RFD-FHEM/SIGNALDuino #22: Extension: Rohrmotor24

thoffma3, 2015-08-06:
Das Protokoll ist entschlüsselt und es gibt schon Eigenbaulösungen. Siehe hier:
http://www.mikrocontroller.net/topic/242784?goto=3962231#3962231

In Github existiert bereits ein Implementierung (aber nicht für Fhem):
https://github.com/bjwelker/Raspi-Rollo

Das ist auch ein gute Quelle 
http://physudo.blogspot.de/2013/08/home-automation-mit-dem-arduino-und-433_17.html
http://physudo.blogspot.de/2013/08/home-automation-mit-dem-arduino-und-433_18.html

sidey79, 2015-08-18:
Das Signal wird doch vom Arduino bereits erkannt oder?

thoffma3, 2015-09-23:
Hallo Sidey. Blicke grad nicht durch. Hab auch einige Wochen nichts gemacht. Hast du das jetzt was implementiert was ich testen kann? Signalduino hab ich grad aktualisiert und ist jetzt bei mir frisch auf dem Fhem

sidey79, 2015-09-23:
Ich gestehe, ich habe mit dem Rohrmotor so gut wie nichts gemacht.
Der Arduino müsste das Signal ja bereits erkennen und an FHEM übergeben. Daher habe ich dein Anliegen referenziert. Der Arduino Code und der FHEM Code liegen in getrennten repos. Ich bau dir mal schnell die Rohrmotor Erkennung ein, dann kannst Du das testen.

Jarnsen, 2015-11-15:
Hallo,
hat sich inzwischen was getan?? Erkannt wird der Code ja. Senden kann ich ihn aber leider nicht. Da passiert nichts. Kann man da was einbauen das es als QuadState gesendet wird?? würde ja erstmal reichen.

sidey79, 2015-11-16:
Hallo Jarnsen,

Leider hatte bislang niemand Signaldaten zur Verfügung gestellt.

Ich kann die Demodulierung jetzt anpassen.
Senden sollte auch gehen. Beispiel kommt später.

Jarnsen, 2015-11-16:
Ok danke...

Wenn ich noch irgendwas beisteuern kann sag bescheid..... Im Microcontroller Forum wurde sogar die ausgelesene Firmenware eines Handsenders hinterlegt...

Stelle mich gern zum testen bereit....

sidey79, 2015-11-16:
details at RFDHEM\RFFHEM#19

closed 2016-09-21 by sidey79