RFD-FHEM/SIGNALDuino #32: zu kleiner Puffer bei Oregon v2 Protokoll wie z.B BTHR918N mit 96 Bit

Ralf9, 2016-02-13:
So wies aussieht ist die Nachrichtenlänge vom BTHR918N mit 96 Bit zu lang.
MC;LL=-1009;LH=947;SL=-534;SH=436;D=AAAAAAAB32CCCB2D555532CD54D4D535554D4D4CB552D2B52A;C=460;
ergibt 
505A6D005A102140448339  dies sind nur 88 Bit
dies ist kurz. Richtig wäre:
605a6d005a1021404483394eef

Bei OSV2 hat die Preamble 32 und der Sync 8 Bit
Beim  BTHR918N  sind es 96 * 2  = 192 Datenbit
Dies ergibt dann 232 Bit

Ralf9, 2016-02-13:
_von Sidey:_

Ich hab noch mal nachgesehen, der Signalpuffer ist 240 Byte. Den muss ich mal auf 254 erhöhen. Ist noch so ein Überbleibsel von früher.

Bei osv2 werden 32 Pulse als Sync übertragen. Bleiben also noch 208.

Bei 96 Bit Nutzdaten macht das 192 Bit im Signal, da alles doppelt übertragen wird.

192 Bit wären im ungünstigsten Fall 382 Pulse. Das passt nicht.
Kann also sein, dass der Puffer voll läuft. Soweit ich mich erinnere, habe ich das abgefangen, aber am Ende müsste ich das debuggen.

Dafür müsste ich mal so viele Daten Senden und prüfen, ob das geht.

Ralf9, 2016-02-13:
Hallo Sidey,

kannst Du mal für Willi eine Testfirmware mit einem auf 254 erhöhten Puffer erstellen?

Ralf9, 2016-02-13:
_von Willi:_

liegt das an der Limitierung des Buffers der Firmware? Wenn ja, könnte man den erhöhen?

Wie ist generell Eurer Algorithmus, wie die Firmware entscheidet, dass die Signale an FHEM über die serielle Schnittstelle geschickt werden. Wartet Ihr bis für eine bestimmte Zeit (Timeout) keine Signale mehr kommen oder ist das davon abhängig, dass der Buffer voll ist?

sidey79, 2016-02-13:
Der Puffer wurde gerade mit dem commit 3918bb9f34c4518b2db9c65a0d5ad7f434b2262b auf 254 erhöht.
Ich habe die Firmware auch im RDF-FHEM\RFFHEM (478004b5aebe0fd2efccb1093367d6add9873262) eingecheckt.

wherzig, 2016-02-13:
Vielen Dank! Ich habe diese Firmware gerade installiert. FHEM zeigt an:
    V 3.2.0-b12 SIGNALduino - compiled at Feb 13 2016 21:34:09
Sollte demnach diese Firmware sein.

Leider ist die Länge immer noch zu gering.

Grüße

Willi
------------
Auszüge:
2016.02.13 22:39:01 4: sduino/msg READ: MC;LL=-1074;LH=945;SL=-528;SH=404;D=55555566599965AAAAA659AA9696A6AAAAA9AA59959A56969998;C=449;
2016.02.13 22:39:01 4: sduino: Found manchester Protocol id 10 clock 449 -> OSV2o3
2016.02.13 22:39:01 4: sduino: OSV2 protocol detected: preamble_pos = 26
2016.02.13 22:39:01 4: sduino: OSV2 protocol converted to hex: (585A6D005A30230084753953) with length (96) bits
2016.02.13 22:39:01 1: OREGON: decoding delay=76 hex=585A6D005A30230084753953
2016.02.13 22:39:01 1: OREGON: msg=585A6D005A30230084753953 bits=88
2016.02.13 22:39:01 4: OREGON: ERROR: Unknown sensor_id=5a6d bits=88 message='585A6D005A30230084753953'.
2016.02.13 22:39:01 4: sduino: Found manchester Protocol id 12 clock 449 -> Hideki protocol
2016.02.13 22:39:01 4: sduino/msg READ: MC;LL=-1067;LH=953;SL=-527;SH=404;D=AAAAAAAACCB332CB55554CB3552D2D4D55555354B32B34AD2D3300;C=452;
2016.02.13 22:39:01 4: sduino: Found manchester Protocol id 10 clock 452 -> OSV2o3
2016.02.13 22:39:01 4: sduino: OSV2 protocol detected: preamble_pos = 33
2016.02.13 22:39:01 4: sduino: OSV2 protocol converted to hex: (585A6D005A30230084753953) with length (96) bits
2016.02.13 22:39:01 4: sduino: Dropped (585A6D005A30230084753953) due to short time or equal msg
2016.02.13 22:39:01 4: sduino: Found manchester Protocol id 12 clock 452 -> Hideki protocol
2016.02.13 22:39:03 4: sduino/msg READ

Ralf9, 2016-02-13:
Hallo Sidey,

ich habe versucht das Nachrichten einlesen und verarbeiten ein wenig zu verstehen.

In der Interruptroutine handleInterrupt() wird die Pulslänge der einzelnen Pulse gemessen und in den FIFO geschrieben.
In der loop() wird der FIFO ausgelesen und an die "SignalDetectorClass decode" übergeben.

In der  SignalDetectorClass::decode wird doDetect()  aufgerufen

In der doDetect() werden dann die Pulse in pattern[] und message[] gespeichert. Wenn die Mindestlänge erreicht ist, dann wird auch verarbeitet.

Mir ist noch nicht klar ob beim manchester Protocol die Nachricht schon während sem einlesen in den manchester speicher geschrieben wird oder ob die Nachricht erst komplett in der message[] gespeichert sein muss.

Sidney, Du darfst gerne diesen Text korrigieren und ergänzen.

sidey79, 2016-02-13:
Die Nachricht wird in den Manchester Decoder übergeben wenn entweder der Puffer (254 Einträge) voll ist, oder wenn der Musterspeicher (6 Einträge) voll ist.
Dann wird geprüft, ob das ganze Manchester konform moduliert wurde.

So viel Signal wie erkannt wurde, geht dann in den Manchester Speicher, der 30 Byte fasst.
Eigentlich sollte aber wieder zurück in den Detector gesprungen werden, wenn das letzte Signal noch ein valides Manchester war und nur wegen Puffer voll ausgelöst wurde. Mag aber sein, dass bei einer ungeraden Anzahl an Pulsen das nicht klappt.

Kannst Du mal den ManchsterDecoder mittels set command deaktivieren?

Ralf9, 2016-02-13:
Hallo Willi,

nun sieht es besser aus.
Ich habe die raw-Nachricht um ein paar Zeichen ergänzt, nun ist das Oregon Modul damit zufrieden.
msg=585A6D005A30230084753953 bits=88
Da die 53 die checksumme ist, müsste ein Workaround im Oregon Modul möglich sein, bis es Sidey dann im nächsten Release 3.3 eingebaut hat.

```
2016.02.13 22:58:05 4: sduinoD/msg get raw: MC;LL=-1074;LH=945;SL=-528;SH=404;D=55555566599965AAAAA659AA9696A6AAAAA9AA59959A569699989898;C=449;
2016.02.13 22:58:05 4: sduinoD: Found manchester Protocol id 10 clock 449 -> OSV2o3
2016.02.13 22:58:05 4: sduinoD: OSV2 protocol detected: bitData_length = 224 preamble_pos = 26
2016.02.13 22:58:05 4: sduinoD: OSV2 protocol converted to hex: (605A6D005A3023008475395311) with length (104) bits
2016.02.13 22:58:05 5: sduinoD: converted Data to (605A6D005A3023008475395311)
2016.02.13 22:58:05 5: sduinoD dispatch 605A6D005A3023008475395311
2016.02.13 22:58:05 5: OREGON: decoding delay=31 hex=605A6D005A3023008475395311
2016.02.13 22:58:05 3: OREGON: Unknown device BTHR918N, please define it
2016.02.13 22:58:05 2: autocreate: define BTHR918N OREGON BTHR918N
..
2016.02.13 22:59:19 4: BTHR918N decoded Oregon: T: 23.3 H: 40 BAT: ok
```

Ralf9, 2016-02-13:
Hallo Willi, 

bitte füge dies mal in die 41_Oregon.pm ein:
```
   OREGON_type_length_key(0x5a6d, 88) =>
   {
    part => 'BTHR918N', checksum => \&OREGON_checksum5, method => \&OREGON_alt_temphydrobaro,
   },
```

sidey79, 2016-02-13:
Hmm, vielleicht lässt sich das Problem ja noch leicht lösen... ich müsste nur mal was zum Testen haben oder halt gute Daten... :)
Im Prinzip wird doch folgendes empfangen: 55555566599965AAAAA659AA9696A6AAAAA9AA59959A569699989898 

Wenn ich zum testen folgendes sende, sollte doch obiges nur empfangen werden, oder?

55555566599965AAAAA659AA9696A6AAAAA9AA59959A569699989898AB

Ralf9, 2016-02-14:
>Die Nachricht wird in den Manchester Decoder übergeben wenn entweder der Puffer (254 Einträge) voll ist, oder wenn der Musterspeicher (6 Einträge) voll ist.
Dann wird geprüft, ob das ganze Manchester konform moduliert wurde.

Hallo Sidey,

Einige Gedanken dazu:

Ist es evtl auch eine Möglichkeit, wenn der Puffer die minlänge (40) erreicht hat auf ein Manchester Signal zu prüfen und dann den Puffer (40 Einträge) im Manchester Speicher zu speichern und dann
entweder so wie die Pulse reinkommen im Manchester Speicher zu speichern, 
oder den Puffer zu löschen und wenn dann wieder die minlänge erreicht ist, wieder den Puffer (40 Einträge) im Manchester Speicher zu speichern.

Kannst Du ganz grob abschätzen, wie lange es ca dauert bis die 40 Einträge verarbeitet und im Manchester Speicher gespeichert sind, 
und wieviele Pulse sind in dieser Zeit ca in den FIFO geschrieben worden?

Geht das schreiben der Pulse in den Manchester Speicher deutlich schneller als wie sie vom Empfänger ankommen? 

Kann Du grob abschätzen wie die Dauer eines Signales mit 100 Pulsen ist. Ich habe da keine Vorstellung.

Gruß Ralf

sidey79, 2016-02-14:
Hallo Ralf,

1. Ein kurzer Puls beim OSV2 dauert um die 500 Mikrosekunden, ein langer 1000. Je nach Bitfolge sind die dann verteilt. Das Signal aus 100 Pulsen muss demnach mindestens 500*100 Mikrosekunden andauern.

2. Es ist grundsätzlich möglich, auch nach 40 Pulsen die Erkennung zu starten. Allerdings ist der Puffer ja extra 254 Einträge groß. Wozu dann jetzt die Erkennung alle 40 Pulse starten? Das macht alles nur wieder etwas komplexer.

3. Das Umwandeln der Signalpulse in Bits geht schneller als das Empfangen. Ohne es gemessen zu haben, nehme ich an, dass wir die Bits nach 1000 Mikrosekunden umgewandelt haben. Was langsam ist, ist die Umwandlung in einen String und die Ausgabe.

4. Den Fifo könnten wir mal auswerten und den Maxwert in eine Variable schreiben. Dann wissen wir es. Ich nehme an, der hat nie mehr als 10 Einträge.

Grüße Sidey

Ralf9, 2016-02-14:
Ok, wenn in der Zeit wo der Puffer mit 254 Einträgen verarbeitet und in den Manchester Speicher geschrieben wird, es sich nicht mehr als ca 20-30 Einträge im FIFO ansammlen, dürfte es passen.

Interessant wäre evtl noch wieviel Einträge es sich im FIFO während der Umwandlung in einen String und der Ausgabe ansammlen.

wherzig, 2016-02-14:
Hallo Ralf,

wenn ich die Größe auf 88 setze, klappt es beim BTHR918N:

2016.02.14 15:27:17 1: OREGON: decoding delay=0 hex=585A6D005A402030447DC55D
2016.02.14 15:27:17 1: OREGON: msg=585A6D005A402030447DC55D bits=88
2016.02.14 15:27:17 1: BTHR918N_5a decoded Oregon: T: 20.4 H: 43 BAT: ok

Grüße

Willi

Ralf9, 2016-02-14:
Hallo Willi,

>Wird allerdings auch häufiger mit nur 80 Bits erkannt.

Dies dürfte daran liegen, daß dies nur ein Workaround ist. Passt der Workaround vorläufig für Dich?
Kommen bei den BTHR918N im Schnitt mindestens alle 5 - 10 Minuten gültige Nachrichten?

Zu welchem Sensor gehört die ID fab8 ?

Bei den MU-Nachrichten sind auch die Störungen, die vom Empfänger emfangen werden, dabei.
Wenn Du keine Sensoren hast die als MU-Nachrichten übertragen werden, kannst Du sie deaktivieren:
`set disableMessagetype unsyncedMU`

Gruß Ral

Ralf9, 2016-02-14:
Ich habe mal im Oregon Modul beim WTGR800 die Länge auf 84 geändert:

```
   # WTGR800 Temp hydro v3
   OREGON_type_length_key(0xfab8, 84) =>
```

Nun passt es:
```
2016.02.14 17:48:39 4: sduinoD/msg get raw: MC;LL=-1051;LH=906;SL=-556;SH=421;D=FFFFD7C74A0D00480184F698C0;C=442;
2016.02.14 17:48:39 4: sduinoD: Found manchester Protocol id 10 clock 442 -> OSV2o3
2016.02.14 17:48:39 4: sduinoD: OSV3 protocol detected: preamble_pos = 18, message_length = 82
2016.02.14 17:48:39 4: sduinoD: OSV3 protocol converted to hex: (54FAB8142C800460C85BC60) with length (84) bits
2016.02.14 17:48:39 5: sduinoD: converted Data to (54FAB8142C800460C85BC60)
2016.02.14 17:48:39 5: sduinoD dispatch 54FAB8142C800460C85BC60
2016.02.14 17:48:39 5: OREGON: decoding delay=0 hex=54FAB8142C800460C85BC60
2016.02.14 17:48:39 5: OREGON: checksum2 = 91 berechnet: 91
2016.02.14 17:48:39 3: OREGON: Unknown device WTGR800_T_1, please define it
2016.02.14 17:48:39 2: autocreate: define WTGR800_T_1 OREGON WTGR800_T_1
..
2016.02.14 17:51:24 4: WTGR800_T_1 decoded Oregon: T: 4.8 H: 86 BAT: ok
```
Gruß Ralf

wherzig, 2016-02-14:
Danke. Läuft erst einmal so. Ich habe derzeit noch alles auf einer Experimentierplatine mit relativ schlechtem Empfang. Werde das ganze dann mal richtig zusammenlöten mit richtigem Antennenstecker/Antenne.

Der FAB( ist der Temperatursensorteil des WRGT800. Die Daten kommen bei SIGNALduino wechselseitig mit 84 und 80 Bits an. Manchmal wird wohl noch eine 0 angehängt. Das passiert beim RFXCOM nicht.

sidey79, 2016-02-14:
Ich habe mir meinen code jetzt noch mal genauer angesehen.

So sollte es sein:

_"Eigentlich sollte aber wieder zurück in den Detector gesprungen werden, wenn das letzte Signal noch ein valides Manchester war und nur wegen Puffer voll ausgelöst wurde."_

Leider ist ee nicht so. Wenn der Puffer voll ist, wird zwar grundsätzlich mit dem Empfang weiter gemacht, aber vorher wird erst mal alles was im Puffer ist seriell ausgegeben...
Da muss ich mir dann dann doch noch mal was überlegen. Ich weiss nur noch nicht genau, wie ich grantiert wieder in die Manchester Ausgabe zurück komme, sollte danach kein Manchester Signal mehr kommen.

wherzig, 2016-02-15:
Hallo,

wenn ich mir den Empfang des X10-Security-Sensors ansehe, bekomme ich folgende Logs.

Kann man damit etwas anfangen.

Grüße

Willi
---------------------

DS10A Kontakt geschlossen

Zu demodulierender Hexcode 29454a847bd700

2016.02.15 18:05:02 4: sduino/msg READ: MU;P0=658;P1=-581;P2=-1720;P4=10748;P5=-2688;D=0101010201020102010102010201020101010102010101020202020102020202010201020202010045010201010102010201020101020102010201010101020101010202020201020202020102010202020100;CP=0;
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 36 -> socket36 matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2016.02.15 18:05:02 4: sduino/msg READ: MU;P0=656;P1=10710;P2=-2730;P3=-586;P4=-1716;D=01203040303030403040304030304030403040303030304030303040404040304040404030403040404030012030403030304030403040303040304030403030303040303030404040403040404040304030404040300;CP=0;
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 36 -> socket36 matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2016.02.15 18:05:02 4: sduino/msg READ: MU;P0=-3556;P1=10708;P2=-2720;P3=550;P4=-583;P5=-1711;D=01234353434343534353435343435343534353434343435343434353535353435353535343534353535343;CP=3;
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 36 -> socket36 matches, trying to demodulate
2016.02.15 18:05:02 4: sduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate

------------------------------------------------------------------
DS10A Kontakt offen

Zu demodulierender Hexcode 29454a04fbd700

2016.02.15 18:06:35 4: sduino/msg READ: MS;P0=546;P1=-587;P2=-1721;P3=-15617;P4=10642;P5=-2802;D=03450102010101020102010201010201020101010101010201010202020202010202020201020102020201;CP=0;SP=3;O;
2016.02.15 18:06:35 4: sduino/msg READ: MU;P0=-2796;P1=544;P2=-587;P3=-1722;P4=-32001;P5=120;D=01213121212131213121312121312131212121212121312121313131313121313131312131213131312145;CP=1;
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 36 -> socket36 matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2016.02.15 18:06:35 4: sduino/msg READ: MU;P0=-1526;P2=10644;P3=-2792;P4=549;P5=-583;D=023454004545454045404540454540454045454545454540454540404040404540404040454045404040454;CP=4;
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 36 -> socket36 matches, trying to demodulate
2016.02.15 18:06:35 4: sduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate

sidey79, 2016-02-15:
Ja, kann man leicht demodulieren, man muss nur im Signalduino das Protokoll definieren.

Passt aber nicht zu diesem Thema oder?

wherzig, 2016-02-15:
Stimmt. habe deshalb einen neues Thema aufgemacht. Ok?

sidey79, 2016-02-19:
Wir müsste das mal testen. Soll ich dir Firmware compilieren, oder klappt das bei euch?

wherzig, 2016-02-22:
Wäre toll, wenn Du die Firmware compilieren könntest. Ich habe bisher noch keine Entwicklungsumgebung für Arduino aufgesetzt. Geht das problemlos unter Ubuntu (ich setze kein Windows ein)?
Könnten die längeren Puffer auch bei X10 helfen?

sidey79, 2016-02-22:
Okay, ist im dev-r32 enthalten.

Für x10 nutzt es leider nichts

closed 2016-09-21 by sidey79