RFD-FHEM/SIGNALDuino #43: unterstützung des CC1101 Empfängers

Ralf9, 2016-12-13:
Hallo,

Hier ist eine neue Version meines test sketches mit den Routinen für den CC1101.
Ich habe inzwischen auch die set- und get Befehle für den CC1101 aus dem 00_cul.pm in die 00_SIGNALduino.pm übernommen.
Wenn es ok ist, werde ich die angepasste 00_SIGNALduino.pm morgen commiten
[cc1101.ino.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/650295/cc1101.ino.txt)

Gruß Ralf

sidey79, 2016-12-13:
Was passiert, wenn der Arduino die Befehle noch nicht kennt. Stürzt er dann aber oder läuft noch alles

Ralf9, 2016-12-13:
Es sind in der get- und set Routine Abfragen drin:
```
  if (($a[1] eq "ccconf" || $a[1] eq "ccreg") && $hash->{version} && $hash->{version} !~ m/cc1101/) {
    return "This command is only available with a cc1101 receiver";
  }
```

```
  if (($cmd eq "freq" || $cmd eq "bWidth" || $cmd eq "rAmpl" || $cmd eq "sens") && $hash->{version} && $hash->{version} !~ m/cc1101/) {
    return "This command is only available with a cc1101 receiver";
  }
```

sidey79, 2016-12-13:
gut, das überblicke ich nicht ganz, wieso Du es aus version holst und nicht anhand der verfügbaren Kommandos realisiert hast

Ralf9, 2016-12-13:
ist so einfacher. Die verfügbaren Kommandos sind nicht im $hash

sidey79, 2016-12-13:
ok, kann ja später noch mal überdacht werden..

Ralf9, 2016-12-13:
ach ja, wenn der Arduino ein Befehl nicht kennt kommt doch sowieso "unsopported Command" zurück

sidey79, 2016-12-14:
Kannst Du auch einen Patch für die aktuelle SIGNALduino Firmware liefern?

Ralf9, 2016-12-14:
nein um dies sauber in die aktuelle SIGNALduino Firmware einbauen sind meine programmier Kenntnisse nicht ausreichend,
Ich müsste dazu versuchen ob ich die SIGNALduino Firmware so anpassen kann, damit ich sie mit der Arduino IDE kompielieren kann.
Ich denke Du hättest gern, daß die CC1101 Routinen in eine eigene Libary ausgelagert werden. Das erstellen einer neuen Libary und class ist mir zu kompliziert.

Ich denke, wenn ich die SIGNALduino Firmware mit der Arduino IDE kompilieren kann und Du mir die neuen CC1101 Libary und class erstellst,  dann müsste ich in der Lage sein es weiter zu programmieren.

Ralf9, 2016-12-15:
Ich versuche gerade die aktuelle SIGNALduino Firmware mit der Arduino IDE zu kompielieren, dabei bekomme ich die folgenden warnings. Ist dies so ok?
```
signalDecoder.cpp:165:62: warning: right shift count >= width of type [enabled by default]
    if (histo[idx2] == 0 || (pattern[idx] ^ pattern[idx2]) >> 31)
                                                              ^
signalDecoder.cpp: In member function 'int8_t SignalDetectorClass::findpatt(int)':
signalDecoder.cpp:576:31: warning: right shift count >= width of type [enabled by default]
   if ((val ^ pattern[idx]) >> 31)
                               ^
signalDecoder.cpp: In member function 'const bool ManchesterpatternDecoder::doDecode()':
signalDecoder.cpp:969:106: warning: right shift count >= width of type [enabled by default]
       if (pClock < maxPulse && (pdec->pattern[pdec->message[l - 1]] ^ pdec->pattern[pdec->message[l]] )>>31)
```

sidey79, 2016-12-15:
Wenn ich gerade mal so nachdenke. Dann ist die Variable eine vom Typ int16. Die hat nur 16 Bit.

Wieso funktioniert da überhaupt etwas
..

sidey79, 2016-12-15:
ich hab mal ein cc1101 namespace und eine eigene .h und cpp Datei angelegt.
Ist noch nicht fertig

Ralf9, 2016-12-15:
ich hab auch etwas weiter 
[RF_Receiver.ino.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/655999/RF_Receiver.ino.txt)
programmiert, da für eine neue library meine Kenntnisse nicht ausreichen, hab ich alles in die RF_Receiver.ino reingepackt

sidey79, 2016-12-16:
Kannst Du deine Ergänzungen in die cc1101.h integrieren?

Eine Lib habe ich nicht erstellt, ich habe das ganze nur in eine cc1101.h ausgelagert und in einen eigenen Namensraum gepackt

Ralf9, 2016-12-16:
Ja, ich werde sie integrieren.

Was ist ein namespace?

sidey79, 2016-12-16:
Ein Namensraum.

In einem Namensraum können Variablen und Funktionen definiert werden, die gleichnamige in einem anderen Namensraum definiert werden.

Um die Funktion in einem Namensraum zu adressieren, kann stellt man den Namen des Namensraumes gefolgt mit :: vor den Funktions oder Variablennamen.

So iat es möglich, die Funktion handleCommand in mehreren Nameapace zu definieren.

Ralf9, 2016-12-16:
Ich habe meine Ergänzungen commitet.
In der checkCC1101() muß noch getested werden ob ein CC1101 angeschlossen wird.
```
  bool checkCC1101() {
    return true;
  }
```

Das Senden funktioniert noch nicht.

Ralf9, 2016-12-30:
Ich habe wegen dem CC1101 Untermenü beim set mal im Forum unter Wunschliste nachgefragt:
https://forum.fhem.de/index.php/topic,63671.msg548927.html#msg548927

Mit FW_detailFn ist es möglich, mir ist das aber zu kompliziert und zu aufwändig.
Mir fehlen die dazu notwendigen html und javascript kenntnisse

sidey79, 2016-12-30:
Danke. Ich habe mir den thread durchgelesen.

Es geht demnach, ist aber Handarbeit.

Ich schau mir das mal an, wenn es zu kompliziert ist, dann überlegen wir uns was anderes.

Grüße Sidey

Ralf9, 2017-01-01:
Ich habe mir mal den html Quelltext vom Auswahlfeld neben dem set angeschaut, mir ist dabei nicht klar wie das funktioniert, daß z.B. beim klicken auf raw ein Eingabefeld und beim klicken auf disableMessagetype ein Auswahlfeld dargestellt wird

```
<div class='makeSelect' dev="sduinoD" cmd="set" list=" close:noArg disableMessagetype:syncedMS,unsyncedMU,manchesterMC enableMessagetype:syncedMS,unsyncedMU,manchesterMC flash raw reset:noArg sendMsg "><form method="post" action="/fhem" autocomplete="off"><input type="hidden" name="detail" value="sduinoD"/><input type="hidden" name="dev.setsduinoD" value="sduinoD"/><input type="submit" name="cmd.setsduinoD" value="set" class="set"/><div class="set downText">&nbsp;sduinoD&nbsp;</div><select  id="sel_setsduinoD" informId="sel_setsduinoD" name="arg.setsduinoD" class="set"><option selected="selected" value='close'>close</option>
<option value='disableMessagetype'>disableMessagetype</option>
<option value='enableMessagetype'>enableMessagetype</option>
<option value='flash'>flash</option>
<option value='raw'>raw</option>
<option value='reset'>reset</option>
<option value='sendMsg'>sendMsg</option>
</select><input type="text" name="val.setsduinoD" class="set" size="30" value=""/></form></div>
```

Ralf9, 2017-01-11:
ist der pro mini 8 MHz, 3,3 V für den signalduino schnell genug?
Ich dachte bei 8 MHz sei die Geschwindigkeit grenzwertig.
Hast Du schon rausgefunden warum beim 8 MHz pro mini die Initialisierung nicht klappt?

sidey79, 2017-01-11:
Das Problem mit dem pro mini liegt nicht am pro mini und auch nicht an den 8 Mhz.

Es liegt eher daran, dass der cc1101 nicht richtig initialisiert ist und der Interrupt dadurch zu oft ausgeführt wird.

Ich habe das verifiziert in dem ich den Interrupt im Quellcode Mal nicht aktiviert hatte.
Dadurch konnte ich die Register mit r99n auslesen.

Die waren alle auf FF gesetzt.

Nach einem factory reset (e) hat es dann funktioniert.

Wir müssten vor dem Starten des Interrupts sicherstellen, dass kompatible Register gesetzt sind. Da scheint noch etwas nicht zu passen.

Bei meinem Nano hatte ich ähnliche Probleme. Der konnte halt die doppelte Anzahl an Instruktionen verarbeiten. Von daher ist das nicht so schwer ins Gewicht gefallen.

Zur Leistungsfähigkeit kann ich jetzt wenig genaues sagen. Ich habe es mal ein paar Minuten laufen lassen und es hat auch mit 8 Mhz geklappt.

Der Flaschenhals ist die Serielle Ausgabe. Sobald der TX Buffer voll ist (64 Byte soweit ich mich erinnere) wird der Arduino blockiert. Das ist die kritische Stelle, welche der FiFo auffangen muss.
Alles andere dauert nur wenige Microsekunden.

Ralf9, 2017-01-11:
> Ich habe das verifiziert in dem ich den Interrupt im Quellcode Mal nicht aktiviert hatte.
> Dadurch konnte ich die Register mit r99n auslesen.
> Die waren alle auf FF gesetzt.

Ich hatte noch nie Probleme beim initialisieren.

Den Befehl r99n für das gesammte EEPROM gibt es nicht. Mit C99 werden alle Register ausgelesen.

Wenn der cc1101 mal wieder nicht richtig initialisiert ist, dann wäre
C99
r00n
r10n
r20n
interessant

sidey79, 2017-01-11:
Hmm, hat man den cc1101 einmal die richtigen Register gesetzt, dann kommt man in das Problem ja nicht mehr rein.

Vielleicht finde ich ja eine Möglichkeit alle Register zu resetten.

Ralf9, 2017-01-15:
kannst Du irgendwie abschätzen oder testen ob der 8 MHz promini für einen stabilen Betrieb schnell genug ist oder ob es grenzwertig ist.
Damit könnte man dann einen CC1101 ohne Levelshifter anschliessen.

sidey79, 2017-01-15:
Man könnte einen Nano und einen pro mini parallel laufen lassen. Ich hatte ja mal kurz einen pro mini laufen. Das sah erst mal normal aus.

Dann würde man sehen, ob es die gleichen Ergebnisse liefert.

Alternativ oder zusätzlich kann man die Zeit der Schleife messen. Die Zeit bis zu einem Nächsten Schleifen Durchlauf muss geringer sein, als das Auslösen des Interrupts. Wenn man das auch noch zum Nano vergleicht, hat man sogar die Differenz.

Schätzen würde ich, dass es in der Regel funktioniert. Bei der cc1101 Variante haben wir ja auch den Vorteil, dass der Interrupt nur auslöst, wenn Daten empfangen wurden. Das ganze Rauschen tritt ja nicht auf.

Der Flaschenhals liegt meiner Meinung nach ohnehin in der seriellen Ausgabe und den String Operationen. Quasi alles was in der Funktion process() läuft.

Ralf9, 2017-01-15:
es dürfte auch wichtig sein, daß die Routinen in der Signaldecoder.ccp die Daten aus dem FIFO schneller abholen als sie die Interruptroutine reinschreibt.
Evtl wäre es auch interessant den freien Platz im FIFO zu vergleichen.
Wenn es nur in der Regel funktioniert ist für einen stabilen Betrieb zu wenig. Es sollte auch noch ausreichend Reserven geben.

Das ist mir nicht klar warum der CC1101 weniger Rauschen empängt, liegt das u.a. an der geringeren Bandbreite?

sidey79, 2017-01-15:
Der cc1101 löst den Interrupt ja nur aus, wenn etwas empfangen wurde.

Beim rxb6 löst der Interrupt auch aus, wenn es nur rauscht.

Da Pegeländerungen kleiner 90 Microsekunden derzeit nicht in den Fifo gelangen sollte das reichen.

Es kann aber sein, dass der Fifo während der seriellen Ausgabe überläuft, da wir mehr senden, als in den TX Puffer passt.

sidey79, 2017-02-02:
Ich habe mal die Initialisierung zwischen der elechohouse lib und unserer Implementierung angesehen

1. Wir haben kürzere delays
2. wir prüfen nicht den Status von MISO_PIN, nicht nach dem SPI Sendebefehl 

```
digitalWrite(SS_PIN, LOW);
	delay(1);
	digitalWrite(SS_PIN, HIGH);
	delay(1);
	digitalWrite(SS_PIN, LOW);
	while(digitalRead(MISO_PIN));
	SpiTransfer(CC1101_SRES);
	while(digitalRead(MISO_PIN));
	digitalWrite(SS_PIN, HIGH);
```

```
cc1101_Deselect();                                  // some deselect and selects to init the cc1101
		delayMicroseconds(5);
		cc1101_Select();
		delayMicroseconds(10);
		cc1101_Deselect();
		delayMicroseconds(41);
		cmdStrobe(CC1101_SRES);                               // send reset
		delay(10);
```

Ralf9, 2017-02-04:
Die delays habe ich aus der askin-lib übernommen, evtl reicht es die delays zu erhöhen.
Hier sind die delays aus der a-culfw: 
```
  my_delay_us(30);                      // Toggle chip select signal
  CC1100_ASSERT;
  my_delay_us(30);
  CC1100_DEASSERT;
  my_delay_us(45);
```

sidey79, 2017-02-04:
soll ich mal auf 30,30 45 erhöhen und eine neue version für den radino compilieren ?

Ralf9, 2017-02-04:
von 5,10,41 auf 30,30,41 ist doch ein erhöhen???

sidey79, 2017-02-04:
ich bin total verwirrt... was unser code war..... ich passe das mal an....

Ralf9, 2017-02-07:
nun ist es klar warum der Signalduino nicht mit dem pro micro funktionieren kann.
Beim ATmega32U4 ist Zuordnung der Arduino Pin zu den Ports ganz anders.
Es gibt nun mehrere Möglichkeiten. Eine Möglichkeit wäre in der output.h die Routinen von der culfw zu verwenden, damit würde es nochmals etwas schneller.
Oder die output.h um den pro mirco zu erweitern 
https://github.com/watterott/Arduino-Libs/blob/master/digitalWriteFast/digitalWriteFast.h

```
#if (defined(ARDUINO_AVR_UNO) || \
       defined(ARDUINO_AVR_DUEMILANOVE) || \
       defined(__AVR_ATmega328__) || \
       defined(__AVR_ATmega328P__) || \
       defined(__AVR_ATmega328PB__))
#define portOfPin(P)\
  (((P)>=0&&(P)<8)?&PORTD:(((P)>7&&(P)<14)?&PORTB:&PORTC))
#define ddrOfPin(P)\
  (((P)>=0&&(P)<8)?&DDRD:(((P)>7&&(P)<14)?&DDRB:&DDRC))
#define pinOfPin(P)\
  (((P)>=0&&(P)<8)?&PIND:(((P)>7&&(P)<14)?&PINB:&PINC))
#define pinIndex(P)((uint8_t)(P>13?P-14:P&7))
#elif (defined(ARDUINO_AVR_LEONARDO) || \
       defined(__AVR_ATmega16U4__) || \
       defined(__AVR_ATmega32U4__))
```

sidey79, 2017-02-07:
Stimmt das ist es.  Arghh sage ich nur.

Ralf9, 2017-03-09:
> Im cc1101 Branch. Der ganz normale C Befehl liefert ja bereits ein abweichendes Ergebnis.

In welcher Zeile hast Du den Code eingefügt?
https://github.com/RFD-FHEM/SIGNALDuino/blob/dev-r33_cc1101/RF_Receiver/RF_Receiver.ino

sidey79, 2017-03-10:
Wir reden irgendwie aneinander vorbei :-(

Das Problem existiert bereits unabhängig einer neuen Funktion.

Hier die Funktion. readReg liefert bei mir nicht das gleiche Ergebnis was in initVal steht.
```
	bool regCheck()
	{
		
		DBG_PRINT("CC1100_PKTCTRL0="); DBG_PRINT(readReg(CC1100_PKTCTRL0, CC1101_CONFIG));
		DBG_PRINT(" vs initVal PKTCTRL0="); DBG_PRINTLN(initVal[CC1100_PKTCTRL0]);

		DBG_PRINT("C1100_IOCFG2="); DBG_PRINT(readReg(CC1100_IOCFG2, CC1101_CONFIG));
		DBG_PRINT(" vs initVal IOCFG2="); DBG_PRINTLN(initVal[CC1100_IOCFG2]);

		return (readReg(CC1100_PKTCTRL0, CC1101_CONFIG) == initVal[CC1100_PKTCTRL0]) && (readReg(CC1100_IOCFG2, CC1101_CONFIG) == initVal[CC1100_IOCFG2]);
	}

```

sidey79, 2017-03-10:
Seltsam...der Code funktioniert heute einwandfrei

Ralf9, 2017-03-10:
ich sehe auch kein Grund warum das readReg nicht funktionieren sollte?
Das auslesen der cc1101 Register mit dem C-Kommando funktioniert ja auch problemlos.

commit 0196d6e7 (2016-12-15): fixed #43 (wrong bitshift for int16)
closed 2019-12-03 by sidey79