RFD-FHEM/SIGNALDuino #48: Same State multiple times in msg

sidey79, 2017-01-27:
`2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=777777777777777777777777777777777;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=77777777777777777777777777777777775;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=77777777777777777777777777777777777;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=7777777777777777777777777777777777775;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=7777777777777777777777777777777777777;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=777777777777777777777777777777777777775;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=777777777777777777777777777777777777777;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=77777777777777777777777777777777777777775;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=77777777777777777777777777777777777777777;CP=-1;SP=3;R=6;
2017.01.27 14:32:27 4: SD433/msg READ: MS;P7=-32001;D=7777777777777777777777777777777777777777775;CP=-1;SP=3;R=6;
2017.01.27 14:32:28 4: SD433/msg READ: MS;P7=-32001;D=7777777777777777777777777777777777777777777;CP=-1;SP=3;R=6;
2017.01.27 14:32:28 4: SD433/msg READ: MS;P7=-32001;D=777777777777777777777777777777777777777777775;CP=-1;SP=3;R=6;

2017.01.27 14:32:46 4: SD433/msg READ: MS;P0=-499;P4=-32001;D=0000000000000000000000000000000000000004444444444;CP=-1;SP=3;R=51;
2017.01.27 14:32:46 4: SD433/msg READ: MS;P0=-499;P4=-32001;D=000000000000000000000000000000000000000444444444442;CP=-1;SP=3;R=51;
2017.01.27 14:32:46 4: SD433/msg READ: MS;P0=-499;P4=-32001;D=000000000000000000000000000000000000000444444444444;CP=-1;SP=3;R=51;
2017.01.27 14:32:46 4: SD433/msg READ: MS;P0=-499;P4=-32001;D=00000000000000000000000000000000000000044444444444441;CP=-1;SP=3;R=51;
`

-> cronjob adds -32001 to fifo
SignalDetectorClass::doDetect  should check for different sign, there is a bug

sidey79, 2017-01-28:
@Ralf9 
Hmm die sache ist sehr seltsam. Ich kann den Fehler nicht finden.

Werte mit gleichem Vorzeichen werden nicht in den Puffer geschrieben und noch besser ist, dass es MS Nachrichten sind.
Hast Du eine Idee, wie man den Fehler provoziert?

Ralf9, 2017-01-28:
Wenn ich mit WS36 (ComandStrobe 36) in den SIDLE State wechsle, bekomme ich ab und zu den folgenden Müll.
Wenn bei "CP=" ein unsinniger Wert steht, sollte die Nachricht verworfen werden

```
MS;P0=334;P1=-32001;P2=109;P3=-1028;P4=-17904;P5=-5440;D=011122132415114211111111111111111111111;CP=-1;SP=4;R=2;
MS;P0=334;P1=-32001;P2=109;P3=-1028;P4=-17904;P5=-5440;D=01112213241511421111111111111111111111112;CP=-1;SP=4;R=2;
MS;P0=334;P1=-32001;P2=109;P3=-1028;P4=-17904;P5=-5440;D=01112213241511421111111111111111111111111;CP=-1;SP=4;R=2;
MS;P0=334;P1=-32001;P2=109;P3=-1028;P4=-17904;P5=-5440;D=0111221324151142111111111111111111111111112;CP=-1;SP=4;R=2;
MS;P0=334;P1=-32001;P2=109;P3=-1028;P4=-17904;P5=-5440;D=0111221324151142111111111111111111111111111;CP=-1;SP=4;R=2;
MS;P0=334;P1=-32001;P2=109;P3=-1028;P4=-17904;P5=-5440;D=011122132415114211111111111111111111111111112;CP=-1;SP=4;R=2;
MS;P0=334;P1=-32001;P2=109;P3=-1028;P4=-17904;P5=-5440;D=011122132415114211111111111111111111111111111;CP=-1;SP=4;R=2;
```

sidey79, 2017-02-05:
Ich schätze ich habe den Fehler gefunden

commit a5863868 (2017-02-05): SignalDetectorClass

call reset, if processMessage was not successfull and pulse curretly processed is not valid (fixes #48)
reset state, when calling getClock, to prevent reporting of last valid clock

cc1101.h
rised delays a little bit (#43)

RF_Receiver.ino
lowered delay 1/2 millisecond after OK from getPing() is transmitted
closed 2019-02-25 by sidey79