RFD-FHEM/SIGNALDuino #49: Inclusion of Came senders possible?

andreasloe, 2017-01-30:
I recently decoded the signal of a (black) Came garage door opener, see http://www.came.com/de/torantriebe-tuerantriebe-schrankensysteme/zubehoer/funkempfaenger.html (they have a simple MC, no rolling code). Do you think it might be possible to add this protocol to the SIGNALDuino? That would be great.

The protocol is as follows. Came sends on 433.92 MHz, AM OOK and has pulses length of 335 us and 670 us. The first pulse is a 335 us LOW. Then, a sequence of HIGHs and LOWs follows with the following microsecs 
` 335 670 (code will be interpreted as 0)` 
`  670 335 (will be interpreted as 1).`
The code itself is then a 12-bit sequence of either 0 or 1 (=4096 possibilities as avertised by Came). The sequence closes with a 15018 footer.

So, for example, a valid code would be for example (I added some space for the {01}-code, see below)
`335   335 670  670 335  670 335  335 670  670 335  670 335  670 335  335 670  670 335  670 335  670 335  335 670    15018`
which would correspond to the code <SyncBit?>011011101110<Footer>.

I hope I made myself clear...

sidey79, 2017-01-30:
Is there signal not already receiver by the microcontroller ans printed in the serial connection?

andreasloe, 2017-01-30:
Unfortunately, I cannot check that at the moment because I am running a CUL (and the tools have not arrived yet). But I hope that the signal can be received by the receiver - I do not know how the arduino will save/report/print these signals then. 

(I recorded them via pilight-raw.)