RFD-FHEM/SIGNALDuino #50: Radino cc1101 support

sidey79, 2017-02-09:


sidey79, 2017-02-09:
@homeAutoUser   ich habe deine Anpassungen etwas abgeändert.

HomeAutoUser, 2017-03-02:
Es scheint aktuell noch einen Bug zu geben.
Die USB "Aussetzer" bei FHEM-Neustart kommen derzeit nur auf dem Gerät in Kombination mir RasPi.
Parallel dazu habe ich noch einen nano, welcher ohne Probleme funktioniert.

Spanish daran, selbiger radinoStick läuft ohne Aussetzer an einem anderen Debian Rechner.

HomeAutoUser, 2017-03-02:
Kann sich jemand erklären worin der Unterschied besteht?

- FHEM läuft, Stick dran, Stick empfängt also DevState initialized | STATE opened
- Stick dran steckend, FHEM Stop, FHEM Start , Stick "ruht" also STATE opened | kein DevState initialized, ernst nachdem ich ein get config Bsp. sende, --> Stick empfängt DevState initialized
- das ganze ist reproduzierbar

Der Nano mit cc1101 läuft sofort.

Timing Problem vielleicht? Ich habe bei meinen Test auch mal den radino mit LED´s nur ein Beispielprogramm gesendet und da war ich auch der Annahme, erst nachdem ich künstliche Pausen einbaute, initialisierte er den CC1101 zuverlässig.

Ideen sind gefragt bzw. Anregungen wie ich dem Phänomen direkter auf den Grund gehen kann.
MfG

sidey79, 2017-03-02:
Kannst Du mal ein Verbose 5 von dem sduino machen und alles was im Log dann auftauch dazu posten wenn er nicht erkannt wird?

HomeAutoUser, 2017-03-02:
Hallo,
Logfileauszug | Geräte Nano + Radino als Device aktiv

> 2017.03.02 21:56:47 0: Featurelevel: 5.8
2017.03.02 21:56:47 0: Server started with 10 defined entities (fhem.pl:13537/2017-02-27 perl:5.020002 os:linux user:fhem pid:2980)
2017.03.02 21:56:48 3: sduino_radino/init: disable receiver (XQ)
2017.03.02 21:56:48 3: sduino_nano_433Mhz/init: disable receiver (XQ)
2017.03.02 21:56:49 3: sduino_radino/init: get version, retry = 0
2017.03.02 21:56:49 2: sduino_radino: initialized. v3.3.1
2017.03.02 21:56:49 3: sduino_radino/init: enable receiver (XE)
2017.03.02 21:56:49 3: sduino_nano_433Mhz/init: get version, retry = 0
2017.03.02 21:56:49 2: sduino_nano_433Mhz: initialized. v3.3.1
2017.03.02 21:56:49 3: sduino_nano_433Mhz/init: enable receiver (XE)
2017.03.02 21:56:51 3: OREGON: Unknown device RTGR328N_17_5, please define it
2017.03.02 21:56:51 2: autocreate: define RTGR328N_17_5 OREGON RTGR328N_17_5
2017.03.02 21:56:51 2: autocreate: define FileLog_RTGR328N_17_5 FileLog ./log/RTGR328N_17_5-%Y.log RTGR328N_17_5
2017.03.02 21:56:51 2: autocreate: define SVG_RTGR328N_17_5 SVG FileLog_RTGR328N_17_5:temp4hum4:CURRENT
2017.03.02 21:56:52 1: Hideki: UNDEFINED sensor 30 detected, code Hideki_30_3
2017.03.02 21:56:52 1: Hideki: UNDEFINED sensor 30 detected, code Hideki_30_3
2017.03.02 21:56:52 2: autocreate: define Hideki_30_3 Hideki Hideki_30_3
2017.03.02 21:56:52 2: autocreate: define FileLog_Hideki_30_3 FileLog ./log/Hideki_30_3-%Y.log Hideki_30_3
2017.03.02 21:56:52 2: autocreate: define SVG_Hideki_30_3 SVG FileLog_Hideki_30_3:temp4hum4:CURRENT
2017.03.02 21:56:53 1: Hideki: UNDEFINED sensor 30 detected, code Hideki_30_1
2017.03.02 21:56:53 1: Hideki: UNDEFINED sensor 30 detected, code Hideki_30_1
2017.03.02 21:56:53 2: autocreate: define Hideki_30_1 Hideki Hideki_30_1
2017.03.02 21:56:53 2: autocreate: define FileLog_Hideki_30_1 FileLog ./log/Hideki_30_1-%Y.log Hideki_30_1
2017.03.02 21:56:53 2: autocreate: define SVG_Hideki_30_1 SVG FileLog_Hideki_30_1:temp4hum4:CURRENT
2017.03.02 21:56:53 2: CUL_TCM97001 Unknown device CUL_TCM97001_169, please define it
2017.03.02 21:56:58 1: Hideki: UNDEFINED sensor 30 detected, code Hideki_30_2
2017.03.02 21:56:58 1: Hideki: UNDEFINED sensor 30 detected, code Hideki_30_2
2017.03.02 21:56:58 2: autocreate: define Hideki_30_2 Hideki Hideki_30_2
2017.03.02 21:56:58 2: autocreate: define FileLog_Hideki_30_2 FileLog ./log/Hideki_30_2-%Y.log Hideki_30_2
2017.03.02 21:56:58 2: autocreate: define SVG_Hideki_30_2 SVG FileLog_Hideki_30_2:temp4hum4:CURRENT
2017.03.02 21:57:04 3: sduino_radino: setting Verbose to: 5
2017.03.02 21:57:10 1: Hideki: UNDEFINED sensor 30 detected, code Hideki_30_4
2017.03.02 21:57:10 4: sduino_radino/msg READ: MU;P0=957;P1=-10344;P3=-1023;P4=304;P5=-523;P6=448;D=010303450365030565630365030365056305636505656365056565656563036565650565636565036565030365656565656565036505630565636565;CP=6;R=39;
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 16 -> Dooya shutter mismatches, aborting
2017.03.02 21:57:10 5: sduino_radino: applying filterfunc SIGNALduino_filterSign
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 21 -> einhell garagedoor matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 21 -> einhell garagedoor mismatches, aborting
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 26 -> remote26 matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 26 -> remote26 mismatches, aborting
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 27 -> remote27 mismatches, aborting
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 28 -> IC Ledspot mismatches, aborting
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: Starting demodulation at Position 11
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 36 -> socket36 matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 36 -> socket36 mismatches, aborting
2017.03.02 21:57:10 5: sduino_radino: applying filterfunc SIGNALduino_compPattern
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: Starting demodulation at Position 2
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 49 -> quigg_gt9000 matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: start pattern for MU Protocol id 49 -> quigg_gt9000 mismatches, aborting
2017.03.02 21:57:10 4: sduino_radino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2017.03.02 21:57:10 5: sduino_radino: Starting demodulation at Position 2
[... cut ...]
2017.03.02 21:59:37 4: sduino_radino/msg READ: MC;LL=-1014;LH=935;SL=-532;SH=448;D=B45AD98C0BC775FE45C;C=488;L=74;R=39;
2017.03.02 21:59:37 4: sduino_radino: Found manchester Protocol id 10 clock 488 -> OSV2o3
2017.03.02 21:59:37 5: sduino_radino: extracted data 0100101110100101001001100111001111110100001110001000101000000001101110100011 (bin)
2017.03.02 21:59:37 4: sduino_radino: Found manchester Protocol id 12 clock 488 -> Hideki protocol
2017.03.02 21:59:37 5: sduino_radino: extracted data 0100101110100101001001100111001111110100001110001000101000000001101110100011 (bin)

HomeAutoUser, 2017-03-02:
Logauszug 2: Gerät radino nur aktiv allein am RasPi

> 2017.03.02 22:02:49 0: Server shutdown
2017.03.02 22:02:49 5: sduino_radino SW: XQ
2017.03.02 22:02:57 1: Including fhem.cfg
2017.03.02 22:02:57 3: telnetPort: port 7072 opened
2017.03.02 22:02:57 3: WEB: port 8083 opened
2017.03.02 22:02:57 3: WEBphone: port 8084 opened
2017.03.02 22:02:57 3: WEBtablet: port 8085 opened
2017.03.02 22:02:57 2: eventTypes: loaded 176 events from ./log/eventTypes.txt
2017.03.02 22:02:57 3: sduino_radino: IDlist MS 0 1 13 14 15 17 2 22 23 25 3 32 33 35 38 4 41 45 6 7
2017.03.02 22:02:57 3: sduino_radino: IDlist MU 16 20 21 24 26 27 28 29 30 31 34 36 37 39 40 42 44 46 48 49 5 50 51 8 9
2017.03.02 22:02:57 3: sduino_radino: IDlist MC 10 11 12 18 43 47
2017.03.02 22:02:57 3: Opening sduino_radino device /dev/serial/by-id/usb-Unknown_radino_CC1101-if00
2017.03.02 22:02:57 3: Setting sduino_radino serial parameters to 38400,8,N,1
2017.03.02 22:02:57 1: sduino_radino/define: /dev/serial/by-id/usb-Unknown_radino_CC1101-if00@38400
2017.03.02 22:02:57 1: sduino_radino/init: /dev/serial/by-id/usb-Unknown_radino_CC1101-if00@38400
2017.03.02 22:02:57 3: sduino_radino device opened
2017.03.02 22:02:57 3: sduino_radino: setting Verbose to: 5
2017.03.02 22:02:57 3: sduino_nano_433Mhz: IDlist MS 0 1 13 14 15 17 2 22 23 25 3 32 33 35 38 4 41 45 6 7
2017.03.02 22:02:57 3: sduino_nano_433Mhz: IDlist MU 16 20 21 24 26 27 28 29 30 31 34 36 37 39 40 42 44 46 48 49 5 50 51 8 9
2017.03.02 22:02:57 3: sduino_nano_433Mhz: IDlist MC 10 11 12 18 43 47
2017.03.02 22:02:57 3: Opening sduino_nano_433Mhz device /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
2017.03.02 22:02:57 3: Can't open /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0: Datei oder Verzeichnis nicht gefunden
2017.03.02 22:02:57 1: Including ./log/fhem.save
2017.03.02 22:02:57 2: SecurityCheck:  WEB,WEBphone,WEBtablet has no associated allowed device with basicAuth. telnetPort has no associated allowed device with password/globalpassword.  Restart FHEM for a new check if the problem is fixed, or set the global attribute motd to none to supress this message. 
2017.03.02 22:02:57 0: Featurelevel: 5.8
2017.03.02 22:02:57 0: Server started with 36 defined entities (fhem.pl:13537/2017-02-27 perl:5.020002 os:linux user:fhem pid:3067)
2017.03.02 22:05:33 5: sduino_radino: command for gets: CG 
2017.03.02 22:05:33 3: sduino_radino/init: disable receiver (XQ)
2017.03.02 22:05:33 5: sduino_radino SW: XQ
2017.03.02 22:05:33 3: sduino_radino/init: get version, retry = 0
2017.03.02 22:05:33 5: sduino_radino SW: V
2017.03.02 22:05:33 5: sduino_radino SW: CG
2017.03.02 22:05:33 4: sduino_radino/msg READ: V 3.3.1-dev Backup_170221 SIGNALduino cc1101 (433Mhz) - compiled at Mar  2 2017 20:17:59
2017.03.02 22:05:33 5: sduino_radino/msg READ: regexp=V\s.*SIGNAL(duino|ESP).* cmd=version msg=V 3.3.1-dev Backup_170221 SIGNALduino cc1101 (433Mhz) - compiled at Mar  2 2017 20:17:59
2017.03.02 22:05:33 2: sduino_radino: initialized. v3.3.1
2017.03.02 22:05:33 5: sduino_radino SW: XE
2017.03.02 22:05:33 3: sduino_radino/init: enable receiver (XE)
2017.03.02 22:05:33 4: sduino_radino/msg READ: MS=1;MU=1;MC=1
2017.03.02 22:05:33 4: sduino_radino/HandleWriteQueue: nothing to send, stopping timer
2017.03.02 22:05:42 4: sduino_radino/msg READ: MC;LL=-1005;LH=953;SL=-509;SH=460;D=A312D16B2E182F1D56B1D7;C=487;L=88;R=24;
2017.03.02 22:05:42 4: sduino_radino: Found manchester Protocol id 10 clock 487 -> OSV2o3
2017.03.02 22:05:42 5: sduino_radino: extracted data 0101110011101101001011101001010011010001111001111101000011100010101010010100111000101000 (bin)
2017.03.02 22:05:42 4: sduino_radino: Found manchester Protocol id 12 clock 487 -> Hideki protocol
2017.03.02 22:05:42 5: sduino_radino: extracted data 0101110011101101001011101001010011010001111001111101000011100010101010010100111000101000 (bin)
2017.03.02 22:05:42 4: sduino_radino/msg READ: MC;LL=-1000;LH=952;SL=-523;SH=455;D=518968B5D70C178EAB5AC58;C=488;L=89;R=24;
2017.03.02 22:05:42 4: sduino_radino: Found manchester Protocol id 10 clock 488 -> OSV2o3
2017.03.02 22:05:42 5: sduino_radino: extracted data 10101110011101101001011101001010001010001111001111101000011100010101010010100101001110100111 (bin)
2017.03.02 22:05:42 4: sduino_radino: Found manchester Protocol id 12 clock 488 -> Hideki protocol
2017.03.02 22:05:42 5: sduino_radino: extracted data 10101110011101101001011101001010001010001111001111101000011100010101010010100101001110100111 (bin)
2017.03.02 22:05:42 4: sduino_radino: hideki protocol converted to hex: 75B7BA8AF1BE3855A52E03 with 92 bits, messagestart 0
2017.03.02 22:05:42 5: sduino_radino: converted Data to (P12#75B7BA8AF1BE3855A52E03)
2017.03.02 22:05:42 5: sduino_radino: dispatch P12#75B7BA8AF1BE3855A52E03
2017.03.02 22:05:42 4: Hideki_Parse sduino_radino incomming P12#75B7BA8AF1BE3855A52E03
2017.03.02 22:05:42 4: Hideki_Parse SensorTyp = 30 decodedString = 75d9ce9e13c248ffef7205
2017.03.02 22:05:42 4: sduino_radino decoded Hideki protocol model=Hideki_30, sensor id=d9, channel=5, temp=21.3, humidity=48, bat=ok, rain=0
2017.03.02 22:05:42 5: deviceCode: Hideki_30_5

sidey79, 2017-03-02:
mach mal ein
`ls -l /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
groups fehm
`

auf der shell vom raspberry pi .

Ich nehme einfach mal an, dein fhem läuft unter dem Benutzer fhem.

HomeAutoUser, 2017-03-02:
root@raspberrypi:/home/pi# ls -l /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 groups fehm
ls: Zugriff auf /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 nicht möglich: Datei oder Verzeichnis nicht gefunden
ls: Zugriff auf groups nicht möglich: Datei oder Verzeichnis nicht gefunden
ls: Zugriff auf fehm nicht möglich: Datei oder Verzeichnis nicht gefunden
root@raspberrypi:/home/pi#

sidey79, 2017-03-02:
Hmm, da gabs einen Fehler im Zeilenumbruch, es handlet sich um zwei Befehle

sidey79, 2017-03-02:
Auf jeden Fall fehlen dem Nutzer root rechte auf /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Ich nehme mal an, das Device gibt es nicht.

HomeAutoUser, 2017-03-02:
Wird vermutlich so sein:
root@raspberrypi:/home/pi# ls -l /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
ls: Zugriff auf /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 nicht möglich: Datei oder Verzeichnis nicht gefunden
root@raspberrypi:/home/pi#

FHEM Benutzer Gruppe nicht vorhanden:
root@raspberrypi:/home/pi# groups fehm
groups: fehm: Einen solchen Benutzer gibt es nicht
root@raspberrypi:/home/pi#

*grübel*
Automatische Installation und dies wurde nicht eingerichtet?

sidey79, 2017-03-02:
Unter welchem Benutzer startest du denn fhem?

Das Problem mit dem Nano ist halt, dass es dieses serielle gerät nicht im Betriebssystem gibt.
Wann genau existiert es denn und wann nicht?

HomeAutoUser, 2017-03-02:
Der Start von FHEM erfolgt mit Sicherheit vom Benutzer pi aus.
pi:x:1000:1000:,,,:/home/pi:/bin/bash
fhem:x:999:20::/opt/fhem:/bin/false

> Wann genau existiert es denn und wann nicht?
Der Fehler oder das Device?
Device /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 existiert immer und so richte ich auch andere Devices ein. Bsp. den Nano.

Fehler: Sobald FHEM gestoppt wird und das Gerät dran steckenbleibt mit einem anschließendem Start FHEM. | FHEM aktiv, Gerät ab und an, dann erkennt er es wieder.

sidey79, 2017-03-02:
Also 
`
Zugriff auf /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 nicht möglich: Datei oder Verzeichnis nicht gefunden
`

Sagt dass es das Device nicht gibt. Dann kann Fhem auch nicht darauf zugreifen. Mir fällt nicht ein, wie FHEM es schaffen soll, dass das Gerät auf der Betriebssysteme ebene verschwindet.
Der Benutzer fhem scheint mir durchaus vorhanden zu sein und die ID 999 zu haben.

HomeAutoUser, 2017-03-03:
Ich habe noch einmal probiert,

> ls -l /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

root@raspberrypi:/home/pi# ls -l /dev/serial/by-id/usb-Unknown_radino_CC1101-if00
lrwxrwxrwx 1 root root 13 Mär  3 13:26 /dev/serial/by-id/usb-Unknown_radino_CC1101-if00 -> ../../ttyACM0
root@raspberrypi:/home/pi#

Also können wir ausschließen, das das Device nicht fehlend ist.
Bei allen Test´s habe ich mitbekommen, das der PING Befehl in FHEM sich anders verhält beim radino als beim NANO. Wie und durch was kommt der PING (nach keinem Meldungsempfang 1Minute) zu stande?

Ich habe die Vermutung, nicht das sogar ein Timing Problem vorliegt, da der radino das USB ja ganz anders "umsetzt" als der NANO.

sidey79, 2017-03-03:
Irgendwie kommunizieren wir aneinander vorbei.

Gestern ging es um einen Nano der nach einem Fhem Neustart nicht mehr gefunden wird. Dazu dazugehörige USB Gerät fehlt schon im Betriebssystem.

Heute geht es um einen radino, der eine andere USB ID hat. Ich dachte, mit dem funktioniert der Fhem Neustart.

Vielleicht kannst Du mir das etwas genauer erklären, was deine Umgebung beinhaltet und mit welchen Komponenten ein Problem besteht.

sidey79, 2017-03-03:
OK hab deinen Beitrag im forum gelesen. Ich denke ich habe das Problem jetzt besser vetstandrn

sidey79, 2017-04-16:
Ich habe das ganze nun versucht nachzustellen.  Mein Aufbau:

raspberrypi1
1x Aruino Nano + angebundenem CC1101
1x ATmega32U4 geflasht als radino ohne angebundenem cc1101

Beide Geräte über /dev/serial-by-id/ in Fhem definiert.
Egal was ich mache, neustarten, fhem stoppen, dann wieder starten etc.

Beide Geräte werden von FHEM erkannt.

HomeAutoUser, 2017-11-04:
Ich denke der Faden kann geschlossen werden.

Der radinoCC1101 funktioniert ggf. bei erneuten Fehlern, einfach reaktivieren.
Bitte Close

closed 2017-11-05 by sidey79