RFD-FHEM/SIGNALDuino #55: Wiederholungen von MS Nachrichten

sidey79, 2017-02-25:
Die Wiederholungen von MS Nachrichten sollen überprüft werden.

Ralf9, 2017-02-26:
Zum herausfinden bei welchen Ids ein überprüfen der Wiederholungen sinnvoll ist, können wir ja ein Attribut z.B. DoubleMsgCheck einführen. In dieses Attribut werden dann Komma getrennt die Ids eingetragen, bei denen nur bei zwei gleichen Nachrichten ($dmsg) in Folge ein Dispatch erfolgt.
Dazu muß noch die $id an die sub SIGNALduno_Dispatch übergeben werden.

```
sub SIGNALduno_Dispatch($$$$$)
{
	my ($hash, $rmsg, $dmsg, $rssi, $id) = @_;
...
```

sidey79, 2017-02-27:
Warum legen wir es nicht einfach in der Protokoll Definition fest? Dann sind alle Konfiguration zum Protokoll an einem Ort.

Ralf9, 2017-02-27:
Ich finde es zum testen und herausfinden bei welchen Ids es sinn macht, praktischer.
Wenn es dann wie gewünscht funktioniert können wir es gerne fest in die Protokoll Definition einbauen.

sidey79, 2017-02-28:
Ich würde Mal sagen, dass es bei allen MS Nachrichten angewendet werden kann.

Ralf9, 2017-02-28:
Ich bin mir nicht sicher ob es auch bei Protokollen mit Prüfsumme sinn macht es anzuwenden.

sidey79, 2017-03-01:
Bei welchen Protokollen haben wir denn einen CRC eingebaut? Mir fällt gerade nichts ein, wo wir das gemacht hätten.

Ralf9, 2017-03-06:
Ja, stimmt das einzige das mir auf Anhieb einfällt ist das 14_CUL_TCM97001, dort haben aber nicht alle Modelle eine checksum.

Ich denke das prüfen der Wiederholungen macht nur bei den Protokollen sinn, bei denen es mindestens zwei Wiederholungen gibt.
Auf Anhieb fallen mir die ProtokollId 7 + 3 ein. Hat das ITv3 auch ausreichend Wiederholungen?

Zum festlegen der IDs für das Prüfen der Wiederholungen fallen mir die folgenden Möglickeiten ein:

- 1  Festlegung in der Protokoll Definition. Per default ist die Prüfung ausgeschaltet. Einschalten über ein Attribut
- 2  Festlegung in der Protokoll Definition. Per default ist die Prüfung eingeschaltet. Aktivierung über ein Attribut
- 3 Festlegung in der Protokoll Definition. Per default ist die Prüfung eingeschaltet. 
Zusätzlich gibt es noch das Attribut z.B. DoubleMsgCheck (s.o.). Wenn es definiert ist, hat es Vorrang vor der Protokoll Definition

Ich würde die dritte Variante bevorzugen.

sidey79, 2017-03-07:
DoublMsgCheck ist ein Fhem Attribut oder ein Signalduino Attribut?
Und wann genau soll es gesetzt werden?

Ralf9, 2017-03-07:
Das DoublMsgCheck ist ein Signalduino Attribut.
Es kann gesetzt werden, wenn die in der Protokoll Definition festgelegten Ids nicht passen. Es kann z.B. definiert werden, wenn bei zusätzlichen IDs eine Wiederholungsprüfung gemacht werden soll.
Oder gehst Du davon aus, daß bei MS-Nachrichten bei allen Protokollen die Wiederholungsprüfung aktiviert werden kann, da es bei allen Ids genügend Wiederholungen gibt?

sidey79, 2017-03-07:
Ja, davon gehe ich aus.

Ich nehmen auch an, dass es zu kompliziert wird.
Wann soll der Anwender das aktivieren/deaktivieren?

Ralf9, 2017-03-07:
In Ausnahmefällen kann es evtl hilfreich sein, wenn es die Möglichkeit gibt, die Wiederholungsprüfung zu deaktivieren. Z.B. wenn der Verdacht besteht, daß bei zuwenig Nachrichten ein dispatch zum Modul gemacht wird.

sidey79, 2017-03-08:
OK, global an und aus wäre für Fehlersuche ein Anwendungsfall.

Das ganze dann noch für einzelne Protokolle oder Sensoren wird wieder zu kompliziert.

closed 2019-02-25 by sidey79