RFD-FHEM/SIGNALDuino #56: Erkenntnis - Erfahrungen DEV CC1101

HomeAutoUser, 2017-06-04:
Hallo @sidey79 

schöne Pfingsten!
Ich habe aktuell deine DEV Version mal getestet und auf den radino gespielt.
Die Version um den 01.05. herum, was deine letzte Version ist, wurde getestet und verglichen mit der Vorgängerversion um den 10.03. herum.

**Die aktuelle Version empfängt 2/3 schlechter als die vorherige. Manche Sensoren empfange ich dadurch sogar gar nicht mehr. Sobald ich diese wieder zurück spiele, so ist der Empfang von den Sensoren abrupt wieder besser und alle Sensoren melden sich**

Diese Erkenntnis wollte ich mal Preis geben. Eine Verbesserung ist in dem Falle noch nicht erzielt, eher eine "Verschlechterung" was den Empfang angeht. Kann man Dir ggf. Zuarbeit leisten um das Verhalten "einzugrenzen"?

Liebe Grüße

sidey79, 2017-06-04:
Hallo.

Ich beobachte teilweise auch, dass es mal schlechter zu funktionieren scheint.
So ganz messen konnte ich das Ganze aber noch nicht wirklich. Wenn ich dem schon nachgehen wollte sah es wieder gut aus.

Bei welchen Protokollen gibt es denn Probleme? Kannst Du mir dazu Signaldaten schicken?

Was das #define ARDUINO_AVR_ICT_BOARDS_ICT_BOARDS_AVR_RADINOCC1101, das ist bei mir automatisch vorhanden, sobald ich das Board auswähle.
Bei dir scheinbar nicht?

HomeAutoUser, 2017-06-13:
Guten Morgen,

> Bei welchen Protokollen gibt es denn Probleme? Kannst Du mir dazu Signaldaten schicken?
Es betrifft unter anderem die Hideki Protokolle zum Beispiel ABER der generelle Empfang ist sehr schlecht. (radino)

MC;LL=-1004;LH=942;SL=-525;SH=456;D=518968B5BC8C178EABE1F38;C=487;L=89;R=0;

Ich habe den Vergleich vollzogen und der Empfang mit selbigen Empfänger nur andere FW brachte 2/3 schlechtere Empfänge bis hin zu keinem Empfang.

> Was das #define ARDUINO_AVR_ICT_BOARDS_ICT_BOARDS_AVR_RADINOCC1101, das ist bei mir automatisch vorhanden, sobald ich das Board auswähle.
Bei dir scheinbar nicht?

Das ist bei mir nicht so, ich durfte dies immer bisher manuell hinzufügen. Ich nutze die Arduino Software. DU?

Desweiteren ist mir aufgefallen, ich habe mehrere Testboards und nun auch eines mit dem CC1101 in Kombination Arduino Nano. Wenn ich die FW vom März cc1101 einspiele empfange ich Daten. Nehme ich die aktuelle FW bzw. die vom Mai, da empfange ich keine Daten. Die Konfiguration PIN´s ist identisch. Hast du eine Idee @sidey79 ? (Nano + CC1101)

Demnächst führe ich einen Test der Empfänger mal parallel durch um da Erkenntnisse noch zu gewinnen.

MfG

HomeAutoUser, 2017-06-14:
Hallo @sidey79 ,

ich habe eine wichtige Erkenntnis zusammen bezüglich des Empfangs gemacht.
Die FW-Versionen bis und einschließlich März liefert "Rohausgaben" im Terminalfenster unter Linux nach folgendem Schema:

> 2017.06.14 10:13:10 4: sduino_USB1/msg READ: MC;LL=-962;LH=999;SL=-462;SH=508;D=95D33C15F3EB52F28;C=488;L=65;

ABER die Version ab Mai bis die jetzige liefern "Rohausgaben" im Terminalfenster wie folgt:

> 2017.06.14 10:13:10 4: sduino_USB0/msg READ: MC;LL=-1009;LH=944;SL=-513;SH=460;D=51B368B5D33C15F3EB52F28;C=487;L=89;R=22;

Unterschied ist ein **fehlendes** Steuerzeichen nach READ, was zur Folge hat, das die Daten nicht verarbeitet werden in FHEM --> kein Empfang. Nimm dies mal bitte unter die Lupe.
Ich habe es hin und her getestet mit den 3 Versionen und dem Testaufbau NANO + CC1101 + ManchesterProtokolle. 

!!! ACHTUNG, im Terminalfenster unter Windows oder via Putty werden die Zeichen unterdrückt, erst die Linuxansicht brachte mich zu dieser Erkenntnis !!!

MfG

sidey79, 2017-06-14:
Das ist ein guter Hinweis.
Ich habe mir den Quellcode angesehen.

Es scheint tatsächlich so, als ob ich die Ausgabe der Steuerzeichen in einen Abschnitt gepackt habe, der nur aktiv ist, wenn die Firmware mit debug Optionen compiliert wird

closed 2017-09-29 by HomeAutoUser