RFD-FHEM/SIGNALDuino #57: Type SD_WS

HomeAutoUser, 2017-08-20:
Hallo,
wisst Ihr wer an diesem Typ arbeitet?
Seit einem Update erhalte ich nun laufende "Phantasie-Sensoren" die absolut nicht plausibel sind.
Da ich im Forum keinen "Direkten" fand, wer dies unter seinen Fittichen hat, so frage ich erstmal hier an, da ich ja die SIGNALDuino FW nutze.

```
SD_WS_33_TH_1 | T: 15.0555555555556 H: 79 -- | --
2017-08-20_08:46:18 SD_WS_33_TH_1 T: 15.0555555555556 H: 79
2017-08-20_08:46:18 SD_WS_33_TH_1 temperature: 15.0555555555556
2017-08-20_08:46:18 SD_WS_33_TH_1 humidity: 79
2017-08-20_08:46:18 SD_WS_33_TH_1 battery: critical
2017-08-20_08:46:18 SD_WS_33_TH_1 channel: 1
2017-08-20_08:46:18 SD_WS_33_TH_1 DMSG: W33#33A0F57D12C
2017-08-20_08:46:18 SD_WS_33_TH_1 RAWMSG: MS;P1=427;P2=-4100;P3=-2086;P4=-8057;D=14131312121313121212131213131313131212121213121312131212121212131213131312131312131212;CP=1;SP=4;R=210;m=2;
2017-08-20_08:46:18 SD_WS_33_TH_1 RSSI: -97
2017-08-20_08:46:18 SD_WS_33_TH_1 absFeuchte: 10.2
2017-08-20_08:46:18 SD_WS_33_TH_1 dewpoint: 11.4

SD_WS_51_TH_12 | T: 106.9 H: 40 -- | --
2017-08-20_08:51:09 SD_WS_51_TH_12 T: 106.9 H: 40
2017-08-20_08:51:09 SD_WS_51_TH_12 temperature: 106.9
2017-08-20_08:51:09 SD_WS_51_TH_12 humidity: 40
2017-08-20_08:51:09 SD_WS_51_TH_12 battery: ok
2017-08-20_08:51:09 SD_WS_51_TH_12 channel: 12
2017-08-20_08:51:09 SD_WS_51_TH_12 trend: rising
2017-08-20_08:51:09 SD_WS_51_TH_12 DMSG: W51#A980C483AC
2017-08-20_08:51:09 SD_WS_51_TH_12 RSSI: -91
2017-08-20_08:51:09 SD_WS_51_TH_12 RAWMSG: MS;P1=542;P2=-9076;P3=-4097;P4=-2061;P6=136;D=1213141314131414131314141414141414131314141413141413141414141413131314131413136314;CP=1;SP=2;R=222;
2017-08-20_08:51:09 SD_WS_51_TH_12 absFeuchte: 297.3
2017-08-20_08:51:09 SD_WS_51_TH_12 dewpoint: 82.3

SD_WS_51_TH_8 | T: 105.1 H: 60 -- | --
2017-08-20_10:13:38 SD_WS_51_TH_8 T: 105.1 H: 60
2017-08-20_10:13:38 SD_WS_51_TH_8 temperature: 105.1
2017-08-20_10:13:38 SD_WS_51_TH_8 humidity: 60
2017-08-20_10:13:38 SD_WS_51_TH_8 battery: ok
2017-08-20_10:13:38 SD_WS_51_TH_8 channel: 8
2017-08-20_10:13:38 SD_WS_51_TH_8 trend: rising
2017-08-20_10:13:38 SD_WS_51_TH_8 RAWMSG: MS;P0=-740;P1=548;P2=-8895;P3=-4095;P4=-2072;P5=-1488;P7=1428;D=1213141314131414131314141414141414131314141414131413141414141314131314131413152170012;CP=1;SP=2;R=229;
2017-08-20_10:13:38 SD_WS_51_TH_8 RSSI: -87.5
2017-08-20_10:13:38 SD_WS_51_TH_8 DMSG: W51#A980C285A8
2017-08-20_10:13:38 SD_WS_51_TH_8 absFeuchte: 420.9
2017-08-20_10:13:38 SD_WS_51_TH_8 dewpoint: 91.1

SD_WS_WH2_0 | T: 150.8 H: 83 -- | --
2017-08-20_00:57:12 SD_WS_WH2_0 T: 150.8 H: 83
2017-08-20_00:57:12 SD_WS_WH2_0 temperature: 150.8
2017-08-20_00:57:12 SD_WS_WH2_0 humidity: 83
2017-08-20_00:57:12 SD_WS_WH2_0 battery: 0
2017-08-20_00:57:12 SD_WS_WH2_0 channel: 0
2017-08-20_00:57:12 SD_WS_WH2_0 RSSI: -94.5
2017-08-20_00:57:12 SD_WS_WH2_0 RAWMSG: MU;P0=-32001;P1=1382;P2=-1441;P3=258;D=0121212123212123212121212323232323232123212321232323232121232121212321232121232323212323232123212321212;CP=3;R=215;
2017-08-20_00:57:12 SD_WS_WH2_0 DMSG: W64#090FD5E453BA8
2017-08-20_00:57:12 SD_WS_WH2_0 absFeuchte: 2126.6
2017-08-20_00:57:12 SD_WS_WH2_0 dewpoint: 144.1
```

Die angelegten Sensoren haben im Logfile auch mehrere Werte und werden somit mehrfach "falsch registriert".

MfG

Ralf9, 2017-08-20:
Die letzten Änderungen waren von @pejonp "WH2 negative Temp Erkennung"
https://github.com/RFD-FHEM/RFFHEM/commit/36dd4fb1765ed95abb20156323979f09f2a1733a

pejonp, 2017-08-20:
die "WH2 negative Temp Erkennung" haben sich aber nur auf das Protocol "64" WH2 bezogen.
Ein log würde uns jetzt hier weiterbringen Verbose=5.

HomeAutoUser, 2017-08-20:
Ein Log wird nachgereicht.
Ich habe soeben den Level auf 5 gestellt und werde nun nach Empfang des "Mülls" dies hier hereinstellen.

-------------

Bei dem Einstellen kam mir eine Idee bzw. Frage auf. Gibt es irgend eine Möglichkeit, das man nur bestimmte Sensoren loggen lassen könnte. Da bei mir nicht nur 5 Sensoren im Einsatz sind, so ist natürlich auch das Logfile reichlich gefüllt bei Level5. Sollte es dies nicht geben, was haltet Ihr ggf. von dieser Möglichkeit welche man mal in Diskussion geben könnte.

sidey79, 2017-08-20:
Aktuell kann man nur die drei Decoder (mu,mc,MS) deaktivieren.

HomeAutoUser, 2017-08-20:
Hallo @pejonp / @sidey79 ,
hierüber das Logfile zu "senden" geht nicht. I
ch habe mir mal 2 Einträge rausgesucht und diese "beschnitten". Da macht sogar Github dicht. Geschweige denn vom kompletten File mit über 400.000 Zeilen mit "nur" 7h.

> You can't comment at this time — your comment is too long (maximum is 65536 characters).

Kann ich das File euch zusenden?

HomeAutoUser, 2017-08-20:
Nach nochmaligen hinsehen, sind die Fehler bisher in den Protokollen

W33#3393397CD8C
W51#9C99CBE6C6
W64#091FD5E482B78

Die zur Verfügung gestellte Datei ist vom Protokoll 33.
Hier der Dateiauszug für den [Fehler P64](https://drive.google.com/open?id=0B8pdBltfE8wkcFJ3bFNSRnJKTW8)

**Anhaltspunkte:**
2017-08-20_15:32:47 SD_WS_WH2_0 T: 150.8
2017-08-20_17:13:24 SD_WS_WH2_0 T: 150.9 H: 69

Datei Fehler P51 wird nachgereicht.

pejonp, 2017-08-21:
@HomeAutoUser 
mach mal bitte ein update W64# sollte nichts mehr anzeigen.

HomeAutoUser, 2017-08-22:
@pejonp 
das Update habe ich vollzogen und lasse es nun testlaufen.

Im anderen [Post](https://github.com/RFD-FHEM/RFFHEM/issues/169) wo ich den PFR-130 ansprach,
da wird dieser nur erkannt mit den Readings battery und mode. Keine Werte.

> Hast du den WH2 Sensoren ?

Persönlich nicht, aber der Nachbar ;-)

pejonp, 2017-08-22:
@homeautouser
Hast du einen pfr-130 ? Wenn ja, was wird angezeigt ?

HomeAutoUser, 2017-08-22:
@pejonp
Ich habe keinen PFR-130. Ich besitze nur einen VENTUS W174.
Dieser arbeitet vermutlich gleich bzw. ähnlich dem PFR-130.

Es  kann durchaus möglich sein, das [deiner ähnlich dem](http://www.tfd.hu/tfdhu/files/wsprotocol/auriol_protocol_v20.pdf) arbeitet.

HomeAutoUser, 2017-08-23:
@pejonp 
Die Fehlinterpretationen des W64# haben sich bisher nicht wieder gemeldet. *Daumen hoch.*

**Weiterhin auf Bugliste:**
Die Fehlinterpretationen des W33# | des W51# sind weiterhin noch vorhanden.
Zusätzlich bei W51# - battery, der _Status crititcal falsch geschrieben_ ist..

HomeAutoUser, 2017-08-24:
Hallo @sidey79 , @Ralf9 , @pejonp 

soeben bin ich auf etwas gestoßen **wo unbedingt nachgebessert werden muss**.
Ein und die selbe BitFolge wird von 2 Modulen ausgewertet. Ein noGo.

> 2017-08-24_10:56:34 SD_WS_33_TH_1 T: 20.2777777777778 H: 65
> 2017-08-24_10:56:34 SD_WS_33_TH_1 temperature: 20.2777777777778
> 2017-08-24_10:56:34 SD_WS_33_TH_1 humidity: 65
> 2017-08-24_10:56:34 SD_WS_33_TH_1 battery: critical
> 2017-08-24_10:56:34 SD_WS_33_TH_1 channel: 1
> 2017-08-24_10:56:34 SD_WS_33_TH_1 RAWMSG: MS;P0=-4162;P1=442;P2=-8076;P3=-2094;D=12131310101313101010131013131313131310131310101310101313131310131013131310131010131313;CP=1;SP=2;R=216;O;
> 2017-08-24_10:56:34 SD_WS_33_TH_1 DMSG: W33#33A04D85160

> 
> 2017-08-24_10:56:34 SD_WS_51_TH_0 T: 28.3 H: 91
> 2017-08-24_10:56:34 SD_WS_51_TH_0 temperature: 28.3
> 2017-08-24_10:56:34 SD_WS_51_TH_0 humidity: 91
> 2017-08-24_10:56:34 SD_WS_51_TH_0 battery: ok
> 2017-08-24_10:56:34 SD_WS_51_TH_0 channel: 0
> 2017-08-24_10:56:34 SD_WS_51_TH_0 trend: rising
> 2017-08-24_10:56:34 SD_WS_51_TH_0 RSSI: -94
> 2017-08-24_10:56:34 SD_WS_51_TH_0 DMSG: W51#9D026C28B0
> 2017-08-24_10:56:34 SD_WS_51_TH_0 RAWMSG: MS;P0=-4162;P1=442;P2=-8076;P3=-2094;D=12131310101313101010131013131313131310131310101310101313131310131013131310131010131313;CP=1;SP=2;R=216;O;

Da bei dem **P33 eine CRC Prüfung vorhanden ist** UND bei **dem P51 nicht**, so würde ich den P51 hinausnehmen erstmal.

MfG

HomeAutoUser, 2017-10-08:
Begülich dem Modul, empfängt **jemand von Euch irgendweinen SD_WS_51** was innerhalb dem Modul der 51 entspricht? 

Ich erhalte diesen dauerhaft mit wechselnden Kanälen wo ich die Channelerkennung vermutlich in Frage stellen würde. Aufzeichnungen und Vergleiche habe ich erst begonnen da es in meinem Falle ein Sensor des Nachbarns ist ;-)

elektron-bbs, 2017-10-08:
Das Problem zeigt sich bei mir neuerdings auch, seit ich den Regenmelder in Betrieb habe. Ändere mal die Zeile für die maximale Länge für das Protokoll 51 in:
			length_max      => '40',
Damit sollte das schon weitestgehend gelöst sein. Die Auswertung dieses Protokolls umfasst sowieso nur 40 Bit.

HomeAutoUser, 2020-01-13:
Das Problem trat nicht mehr auf und sollte es wieder auftreten, bitte einen Issues hier https://github.com/RFD-FHEM/RFFHEM/issues weil es sich mit hoher Warscheinlichkeit um ein Softwareproblem handelt. Kein Firmware Problem.

closed 2020-01-13 by HomeAutoUser