RFD-FHEM/SIGNALDuino #60: MS | MU Rohdatenausgabe mit Steuerzeichen ab > dev170310 - Bug?

HomeAutoUser, 2017-08-25:
Hallo,
kannst du Dir das erklären @sidey79?
Ich bin bei einer Fehlersuche an die Rohdaten gegangen via Putty um erstmal einiges zu kontrollieren.
(FHEM Schnittstelle ist geschlossen)
Dabei fiel mir auf, das ich bei der dev170310 eine klare leserliche Ausgabe erhalte. Bei sämtlichen Versionen danach, bis zur devmc170822 sind die **MS und MU Ausgaben** mit Steuerzeichen "verstümmelt". 

**DEV170310**
> =~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2017.08.25 09:41:40 =~=~=~=~=~=~=~=~=~=~=~=
Reading values fom eeprom
CCVersion=20
CCPartnum=0
CC1101 found
Starting timerjob
receiver enabled
MS;P1=-4052;P2=487;P3=-9077;P4=-2014;D=23212124212424242124212124212124242421212424242424242424242424242421242421;CP=2;SP=3;R=73;O;
MS;P0=474;P1=-2025;P2=-4049;P3=-9074;D=03020201020101010201020201020201010102020101010101010101010101010102010102;CP=0;SP=3;R=71;O;
MS;P0=476;P1=-2019;P2=-4061;P3=-9080;D=03020201020101010201020201020201010102020101010101010101010101010102010102;CP=0;SP=3;R=71;
MS;P1=-4048;P2=475;P3=-9075;P4=-2021;D=23212124212424242124212124212124242421212424242424242424242424242421242421;CP=2;SP=3;R=71;O;
MS;P0=476;P1=-2022;P2=-4062;P3=-9065;D=03020201020101010201020201020201010102020101010101010101010101010102010102;CP=0;SP=3;R=71;O;
MS;P0=468;P1=-2022;P2=-4054;P3=-9082;D=03020201020101010201020201020201010102020101010101010101010101010102010102;CP=0;SP=3;R=72;

**DEV170615**
>=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2017.08.25 09:36:31 =~=~=~=~=~=~=~=~=~=~=~=
Reading values fom eeprom
CCInit SRES Started
POR Done
CCVersion=20
CCPartnum=0
CC1101 found
Starting timerjob
CC1100_PKTCTRL0=50 vs EEPROM PKTCTRL0=50
C1100_IOCFG2=13 vs EEPROM IOCFG2=13
receiver enabled
Ms;±Ý;’Ú;£ë£;´ë‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;m2;
Ms;±Ð;’é;£ë£;´Ú‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;m1;
Ms;±Ó;’Ý;£í£;´ì‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;m0;
Ms;±Ø;’Ü;£â£;´ä‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;
Mu;ß;±ã‡;²Í;£ö£;D@;C0;R49;
Ms;±Õ;’â;£þ£;´é‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;m2;
Ms;±×;’Ò;³£;´ñ‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;m1;
Ms;±×;’Ü;£ö£;´å‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;m0;
Ms;±Ö;’ß;£ó£;´ß‡;D#!!$!$$$!$!!$!!$$$!!$$$$$$$$$$$$$!$$!;C2;S3;R49;O;
Mu;Ò;±ã‡;²Ú;£ø£;D@;C0;R48;
Mu;€’‚;±¿€;’ø€;³À;„ý;•­€;d5DDDDDDDDDDDDD;C4;R48;

**DEV170822**
> =~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2017.08.25 10:21:03 =~=~=~=~=~=~=~=~=~=~=~=
Reading values fom eeprom
CCInit SRES Started
POR Done
CCVersion=20
CCPartnum=0
CC1101 found
Starting timerjob
CC1100_PKTCTRL0=50 vs EEPROM PKTCTRL0=50
C1100_IOCFG2=13 vs EEPROM IOCFG2=13
receiver enabled
Ms;°ñ‡;‘Ö;£ÿ£;´Ù;d1AAAAAAAA;C1;S3;R45;O;m2;
Ms;°î‡;‘Ï;³ƒ£;´Û;D;C1;S3;R45;O;m1;
Ms;°ï‡;‘Ó;³ƒ£;´á;D;C1;S3;R45;O;m0;
Ms;°î‡;‘Ð;³£;´Û;D;C1;S3;R45;O;
Mu;Ð;±é‡;²á;³„£;D@;C0;R47;
Ms;‘Ù;²î‡;¤ï£;µÔ;d!A!!!!!!!Q!QQ!QQ!!Q!!!!!!!!!!!!!!!Q!!Q;C1;S4;R44;O;m2;
Ms;‘Î;²î‡;¤÷£;µç;D;C1;S4;R44;O;m1;
Ms;‘Ó;²ï‡;¤û£;µâ;D;C1;S4;R44;O;m0;
Ms;‘Ñ;²ó‡;¤ÿ£;µÝ;D;C1;S4;R44;O;
Mu;Ô;±ó‡;²Û;£è£;D@;C0;R45;

In FHEM werden die Daten aber leserlich angezeigt und ggf. verarbeitet.

Ich habe es mit einem Arduino NANO + CC1101_433 bzw. CC1101_868 getestet.
Einstellungen in Putty wurden nicht geändert bzw. diese wurden auch mehrmals verglichen. 
Der einzige Unterschied ist wirklich nur die Firmware. Ältere drauf = lesbar, neuere Drauf, nicht lesbar. Vorfall ist reproduzierbar.

Es wäre schön, wenn wir darauf eine Antwort finden und man die Daten wieder leserlich entnehmen könnte.

MfG

sidey79, 2017-08-25:
Works as desgned.  
Die Datenausgabe wird komprimiert und für uns menschen sieht es wie Steuerzeichen aus.

@Ralf9 
Einen schalter zum aktivieren hatten wir nicht eingebaut oder? Konnte auf Anhieb gerade keinen austindig machen.

Ralf9, 2017-08-25:
Ohne die komprimierung konnte es vorher vorkommen, daß der Empfangspuffer ab und zu übergelaufen ist, da die Daten nicht schnell genug verarbeitet und seriell übertragen werden konnten.
Beim 3,3V promini kamen die Empfangspuffer überläufe häufiger vor, da er nur 8 MHz hat.
Durch die Datenkomprimierung kann der Empfangspuffer auf 100 erhöht werden.

Ja, im git ist noch kein Schalter eingebaut.
Der Schalter lässt sich z.B. in die config einbauen
CER = ein
CDR = aus
`
config   MS=1;MU=1;MC=1;Mred=1;`

u.a.
```
void storeFunctions(const int8_t ms, int8_t mu, int8_t mc, int8_t red)
{
	mu=mu<<1;
	mc=mc<<2;
	red=red<<3;
	int8_t dat =  ms | mu | mc | red;
    EEPROM.write(addr_features,dat);
}

void getFunctions(bool *ms,bool *mu,bool *mc, bool *red)
{
    int8_t dat = EEPROM.read(addr_features);

    *ms=bool (dat &(1<<0));
    *mu=bool (dat &(1<<1));
    *mc=bool (dat &(1<<2));
    *red=bool (dat &(1<<3));
}

inline void configCMD()
{
  bool *bptr;

  if (cmdstring.charAt(2) == 'S') {  	  //MS
	bptr=&musterDec.MSenabled;;
  }
  else if (cmdstring.charAt(2) == 'U') {  //MU
	bptr=&musterDec.MUenabled;;
  }
  else if (cmdstring.charAt(2) == 'C') {  //MC
	bptr=&musterDec.MCenabled;;
  }
  else if (cmdstring.charAt(2) == 'R') {  //Mreduce
	bptr=&musterDec.MredEnabled;;
  }

  if (cmdstring.charAt(1) == 'E') {   // Enable
	*bptr=true;
  }
  else if (cmdstring.charAt(1) == 'D') {  // Disable
	*bptr=false;
  } else {
	return;
  }
  storeFunctions(musterDec.MSenabled, musterDec.MUenabled, musterDec.MCenabled, musterDec.MredEnabled);
}
```

HomeAutoUser, 2017-08-27:
Hallo,
vielen Dank für die Information.
Es wäre schön wenn ggf. der Schalter eingebaut wird.

@Ralf, mit Empfangspuffer überlaufen meint Ihr wohl "Add Dataoverflow" oder wie sich das nennt?

Dataoverflow erscheint bei dem zusammengebauten 868Mhz bei mir, sobald ich einen bestimmten Sender drücke und der als MS oder MC bzw MU "decodiert" wird. auf den ersten Blick sah dies aus als ob der CC1101 sich da nicht entscheiden kann.

Ralf9, 2017-08-27:
Da ist anscheinend noch ein Fehler im Code drin. Das "addData overflow!!" dürfte normal nie kommen.
Ich hatte das "addData overflow!!" auch schon mal.
Ich hatte schon mal versucht den Fehler einzugrenzen, bei mir kommt das "addData overflow!!" nur extrem selten.
Ich habe in der signalDecoder.cpp beim "addData overflow!!" zusätzliche Ausgaben eingebaut.
Vielleicht kann mit diesen zusätzlichen Ausgaben Sidey den Fehler eingrenzen.
https://github.com/RFD-FHEM/SIGNALDuino/commit/1484b4636646afa4802cdae1e43734b684010b73

Ich weiß nicht ob so wie ich den Schalter bei mir eingebaut habe für Sidey ok ist oder ob er eine andere Idee hat wie man den Schalter einbauen kann.

HomeAutoUser, 2017-08-28:
Hallo,
zum Thema addData overflow!!,
bei der aktuellen dev-r33-fixmc erhalte ich nun sporadisch auch erneut addData overflow!!

> 2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!
2017.08.28 17:35:25 4: Arduino_Nano/msg READ: addData overflow!!

Vielleicht hängt es mit Zeiten zusammen? Bei mir wimmelt es ja von Sensoren und da ist die FW nur am arbeiten.

HomeAutoUser, 2017-08-28:
Hallo @Ralf9 
hier sind schon mal Erkenntnisse.

> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=6 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!

closed 2018-04-05 by HomeAutoUser