RFD-FHEM/SIGNALDuino #62: addData overflow

Ralf9, 2017-08-28:
In der aktuellen Version mit message kompression kann es unter bestimmten speziellen recht seltenen Umständen zu einem "addData overflow" kommen.
Vielleicht lässt sich mit diesen zusätzlichen Ausgaben der Fehler eingrenzen:
https://github.com/RFD-FHEM/SIGNALDuino/commit/1484b4636646afa4802cdae1e43734b684010b73

siehe auch
https://github.com/RFD-FHEM/SIGNALDuino/issues/60#issuecomment-325389952

HomeAutoUser, 2017-08-28:
Sorry, falsches Fenster. Daher nun im Richtigen Faden.

Hallo @Ralf9
hier sind schon mal Erkenntnisse.

> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=6 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
> 2017.08.28 19:20:52 4: Arduino_Nano/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!

Ralf9, 2017-08-28:
@sidey79 
reichen Dir diese Ausgaben zur Fehlereingrenzung oder benötigst Du in der Ausgabe weitere Variablen?

HomeAutoUser, 2017-08-31:
Hallo,
ich habe soeben einen "ähnlichen" Data overflow Effekt oder es ist dieser aber das ich nun nicht mehr den Empfänger zur Ausgabe überreden kann.

> 2017.08.31 20:49:52 3: Arduino_Nano: setting Verbose to: 5
2017.08.31 20:50:02 3: Arduino_Nano reset
2017.08.31 20:50:02 3: Opening Arduino_Nano device /dev/ttyUSB0
2017.08.31 20:50:02 3: Setting Arduino_Nano serial parameters to 57600,8,N,1
2017.08.31 20:50:02 1: Arduino_Nano/define: /dev/ttyUSB0@57600
2017.08.31 20:50:02 1: Arduino_Nano/init: /dev/ttyUSB0@57600
2017.08.31 20:50:02 3: Arduino_Nano device opened
2017.08.31 20:50:04 3: Arduino_Nano/init: disable receiver (XQ)
2017.08.31 20:50:04 5: Arduino_Nano SW: XQ
2017.08.31 20:50:04 3: Arduino_Nano/init: get version, retry = 0
2017.08.31 20:50:04 5: Arduino_Nano SW: V
2017.08.31 20:50:14 3: Arduino_Nano/init: get version, retry = 1
2017.08.31 20:50:14 5: Arduino_Nano SW: V
2017.08.31 20:50:24 3: Arduino_Nano/init: get version, retry = 2
2017.08.31 20:50:25 5: Arduino_Nano SW: V
2017.08.31 20:50:35 3: Arduino_Nano/init: get version, retry = 3
2017.08.31 20:50:35 2: Arduino_Nano/init retry count reached. Reset
2017.08.31 20:50:35 3: Arduino_Nano reset
2017.08.31 20:50:35 3: Opening Arduino_Nano device /dev/ttyUSB0
2017.08.31 20:50:35 3: Setting Arduino_Nano serial parameters to 57600,8,N,1
2017.08.31 20:50:35 1: Arduino_Nano/define: /dev/ttyUSB0@57600
2017.08.31 20:50:35 1: Arduino_Nano/init: /dev/ttyUSB0@57600
2017.08.31 20:50:35 3: Arduino_Nano device opened
2017.08.31 20:50:36 3: Arduino_Nano/init: disable receiver (XQ)
2017.08.31 20:50:36 5: Arduino_Nano SW: XQ
2017.08.31 20:50:37 3: Arduino_Nano/init: get version, retry = 0
2017.08.31 20:50:37 5: Arduino_Nano SW: V
2017.08.31 20:50:47 3: Arduino_Nano/init: get version, retry = 1
2017.08.31 20:50:47 5: Arduino_Nano SW: V
2017.08.31 20:50:57 3: Arduino_Nano/init: get version, retry = 2
2017.08.31 20:50:57 5: Arduino_Nano SW: V
2017.08.31 20:51:07 3: Arduino_Nano/init: get version, retry = 3
2017.08.31 20:51:07 2: Arduino_Nano/init retry count reached. Closed
2017.08.31 20:51:07 2: Arduino_Nano closed

Sowas ähnliches konnte ich mal reproduzieren als ich nur auf der Schnittstelle ohne FHEM mir die Daten ansah. Irgendwann stürzte der Empfänger ab und sendete immer eine V.

Firmware derzeit, aktuelle DEV-r33_cc1101 vom 30.08.2017.
Empfänger abstecken und komplett neu starten, dann geht es wieder.

Grüße

HomeAutoUser, 2017-09-05:
Hallo,

hier mal eine andere Konstellationen der Ausgaben.

> 2017.09.05 00:03:53 4: Arduino_Nano/msg READ: val=7 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!
2017.09.05 00:03:53 4: Arduino_Nano/msg READ: val=7 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!
2017.09.05 00:03:53 4: Arduino_Nano/msg READ: val=7 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!
2017.09.05 00:03:53 4: Arduino_Nano/msg READ: val=7 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!

HomeAutoUser, 2017-09-16:
Hallo,

ich habe mir nochmal bestmöglich die Zusammenhänge angesehen oder probiert wann der Fehler auftritt.

Meine Erkenntnisse sind wie folgt:
- **Auftreten des add Data overflow! nur bei AKTIVEN MC-Protokoll**
(verglichen über Tage und sobald ich MC aktiviere, so folgt der Fehler)
- sämtliche Einträge im Logfile welche mit MC zu tun haben, besitzen zusätliche Steuerzeichen

Eine Zusammenstellung des Logfile, wo MC Steuerzeichen sichtbar sind und bei sämtlichen anderen Nachrichten nicht: (geschriebenes Logfile mit NotePad++ geladen und angesehen)

> 2017.09.16 11:12:23 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-19020;P1=396;P2=-394;P3=598;P4=-595;D=012121212121212121212121234121212343412123434123412343412343434121234341212121212341212121212341212121234341212341234;CP=1;R=250;
2017.09.16 11:12:33 4: Arduino_Nano_868Mhz/msg READ: C0Dn11=2165E857C43023B900070018146C070091
2017.09.16 11:12:33 5: Arduino_Nano_868Mhz/msg READ: regexp=C0Dn11.* cmd=ccconf msg=C0Dn11=2165E857C43023B900070018146C070091
2017.09.16 11:12:41 4: Arduino_Nano_868Mhz/msg READ: V 3.3.1-dev_170827 SIGNALduino cc1101 - compiled at Aug 27 2017 22:30:22
2017.09.16 11:12:41 5: Arduino_Nano_868Mhz/msg READ: regexp=V\s.*SIGNAL(duino|ESP).* cmd=version msg=V 3.3.1-dev_170827 SIGNALduino cc1101 - compiled at Aug 27 2017 22:30:22
2017.09.16 11:12:44 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1008;LH=946;SL=-527;SH=443;D=51D668B59C7C1732AB7E938;C=487;L=89;R=220;
2017.09.16 11:12:44 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1006;LH=946;SL=-523;SH=452;D=A8EB345AEE3E0B9955BE5EC;C=487;L=90;R=220;
2017.09.16 11:12:44 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-995;LH=951;SL=-531;SH=462;D=51D6688;C=489;L=26;R=220;
2017.09.16 11:12:44 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1000;LH=943;SL=-511;SH=462;D=ADE3E0B9955BFC4C;C=485;L=62;R=219;
2017.09.16 11:12:45 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.09.16 11:12:47 5: Arduino_Nano_868Mhz: command for gets: CG 
2017.09.16 11:12:47 5: AddSendQueue: Arduino_Nano_868Mhz: CG (1)
2017.09.16 11:12:47 5: Arduino_Nano_868Mhz SW: CG
2017.09.16 11:12:47 4: Arduino_Nano_868Mhz/msg READ: MS=1;MU=1;MC=1
2017.09.16 11:12:47 5: Arduino_Nano_868Mhz/msg READ: regexp=^MS.*MU.*MC.* cmd=config msg=MS=1;MU=1;MC=1
2017.09.16 11:12:47 4: Arduino_Nano_868Mhz/HandleWriteQueue: nothing to send, stopping timer
2017.09.16 11:13:00 4: Arduino_Nano_868Mhz/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:13:01 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:13:01 4: Arduino_Nano_868Mhz/msg READ: val=2 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:13:01 4: Arduino_Nano_868Mhz/msg READ: val=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:13:01 4: Arduino_Nano_868Mhz/msg READ: val=4 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!

> 2017.09.16 11:23:49 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-610;LH=617;SL=-404;SH=392;D=FFFBDADED5D557AEABDDF5F8;C=337;L=93;R=2;
2017.09.16 11:23:49 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=408;P2=-400;P5=653;P6=-582;D=012121212121212121212121256121212561256561256121212561256565612125656565656561212125656561256565612121212561212561212125612121212121256;CP=1;R=72;
2017.09.16 11:23:49 4: SIGNALduino_unknown incomming msg: u72#168B9F8EE12204
2017.09.16 11:23:49 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=398;P2=-410;P5=586;P6=-600;D=012121212121212121212121256121212561256561256121212561256565612125656565656561212125656561256565612121212561212561212125612121212121256;CP=1;R=73;
2017.09.16 11:23:59 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1018;LH=941;SL=-520;SH=448;D=A8EB345ACBC60B9955E0F04;C=487;L=90;R=218;
2017.09.16 11:23:59 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-998;LH=942;SL=-532;SH=450;D=51D668B5D78C1732ABC3CE8;C=486;L=89;R=218;
2017.09.16 11:23:59 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1005;LH=944;SL=-521;SH=455;D=A8EB345ADBC60B9955E07D4;C=487;L=90;R=219;
2017.09.16 11:24:15 4: Arduino_Nano_868Mhz/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:24:17 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:24:17 4: Arduino_Nano_868Mhz/msg READ: val=6 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:24:22 4: Arduino_Nano_868Mhz/msg READ: val=2 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:24:22 4: Arduino_Nano_868Mhz/msg READ: val=0 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.16 11:24:22 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!

> 2017.09.16 11:24:44 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1008;LH=939;SL=-518;SH=452;D=51D668B59C7C1732AB7E938;C=486;L=89;R=218;
2017.09.16 11:24:44 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1008;LH=934;SL=-530;SH=449;D=51D668B5DC7C1732AB7CBD8;C=486;L=89;R=223;
2017.09.16 11:24:44 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1011;LH=938;SL=-523;SH=458;D=A8EB345ADE3E0B9955BFC4C;C=488;L=90;R=219;

> 2017.09.16 11:34:45 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-608;LH=592;SL=-401;SH=376;D=FFFAAD5576DEAADFDAB7D;C=329;L=84;R=38;
[... cut ...]
2017.09.16 11:41:02 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!
2017.09.16 11:41:02 4: Arduino_Nano_868Mhz/msg READ: val=2 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!
2017.09.16 11:41:02 4: Arduino_Nano_868Mhz/msg READ: val=3 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!
2017.09.16 11:41:02 4: Arduino_Nano_868Mhz/msg READ: val=4 msglen=253 bytecnt=126 valcnt=253 mTrunc=1 addData overflow!!

@sidey79 / @Ralf9 könnt Ihr das vielleicht bestätigen oder auch validieren.

MfG

Ralf9, 2017-09-16:
mTrunc=1 bedeuted, daß der buffer voll war und es ein bufferMove gegeben hat. Beim bufferMove müsste normalerweise auch die messageLen verringert werden. Da scheint irgendwas nicht zu passen.

Evtl bekommen wir neue Erkenntnisse, wenn Du am Anfang der 
`void SignalDetectorClass::bufferMove(const uint8_t start)`
die folgende Ausgabe einfügst:
```
MSG_PRINT("start="); MSG_PRINT(start);
MSG_PRINT(" msglen="); MSG_PRINT(messageLen);
MSG_PRINTLN(F(" bufferMove!"));
```

> Bei 5 Versuchen, bekomme ich 4x add Dataoverflow

Bei diesem häufigen auftreten kannst Du auch mal die debug Ausgabe aktivieren.
Dazu mußt Du in der "signalDecoder.h" folgendes einfügen:
#define DEBUGDETECT 2
#define DEBUGDECODE 2

HomeAutoUser, 2017-09-17:
Ich habe nun mal 

> MSG_PRINT("start="); MSG_PRINT(start);
> MSG_PRINT(" msglen="); MSG_PRINT(messageLen);
> MSG_PRINTLN(F(" bufferMove!"));

eingebaut und schaue mal was das System sagt. Eines ist auf jedenfall schon erkennbar, das ein bufferMove vorhanden ist und die msglen sich immer verringert.

> 2017.09.17 14:42:00 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1006;LH=946;SL=-514;SH=459;D=8F3EB0D546C0A0;C=487;L=54;R=254;
2017.09.17 14:42:00 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=11 bufferMove!
2017.09.17 14:42:00 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1053;LH=940;SL=-526;SH=451;D=182688;C=494;L=21;R=254;
2017.09.17 14:42:00 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1019;LH=920;SL=-526;SH=452;D=D6DC3053CAAE4307;C=486;L=64;R=254;
2017.09.17 14:42:03 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1021;LH=929;SL=-523;SH=452;D=5CED2E94C707D5E2A960CE0;C=487;L=89;R=252;
2017.09.17 14:42:03 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1013;LH=941;SL=-507;SH=462;D=2D16BB8F82A1D569B6D8;C=487;L=77;R=252;
2017.09.17 14:42:03 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1009;LH=933;SL=-526;SH=463;D=5CED2E948707D5E2A962FA0;C=488;L=89;R=253;
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=12 msglen=14 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=10 msglen=12 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=3 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=5 msglen=8 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=14 msglen=17 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=4 msglen=6 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=6 msglen=6 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=13 msglen=13 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=16 msglen=16 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=11 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=12 msglen=12 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=16 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=7 msglen=8 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=5 msglen=5 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=11 msglen=12 bufferMove!
2017.09.17 14:42:04 4: Arduino_Nano_868Mhz/msg READ: start=11 msglen=11 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=4 msglen=16 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=5 msglen=15 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=10 msglen=13 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=7 msglen=9 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=7 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=13 msglen=16 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=4 msglen=5 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=5 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=10 msglen=13 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=10 msglen=17 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=16 msglen=19 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=9 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=10 msglen=10 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=20 msglen=21 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=12 msglen=12 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=11 msglen=14 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=9 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=10 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=5 msglen=9 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=8 msglen=12 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=7 msglen=10 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=10 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=9 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=12 msglen=14 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=5 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=7 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=13 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=4 msglen=7 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=8 msglen=9 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=13 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=6 msglen=6 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=8 msglen=11 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=12 msglen=15 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=4 msglen=7 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=6 msglen=9 bufferMove!
2017.09.17 14:42:05 4: Arduino_Nano_868Mhz/msg READ: start=4 msglen=5 bufferMove!
[... cut ...]
2017.09.17 14:42:12 4: Arduino_Nano_868Mhz/msg READ: Ms;��;�ď;�ڇ;��;D%##$#$$$#$##$##$$$#$#$$$$$$$$$$$$#$##;C2;S5;R3A;O;start=74 msglen=254 bufferMove!
2017.09.17 14:42:12 4: Arduino_Nano_868Mhz/msg READ: m1;
2017.09.17 14:42:13 4: Arduino_Nano_868Mhz/msg READ: Ms;�Ձ;�Ώ;��;�ܣ;D%##$#$$$#$##$##$$$#$#$$$$$$$$$$$$#$##;C2;S5;R3A;O;start=74 msglen=254 bufferMove!
2017.09.17 14:42:13 4: Arduino_Nano_868Mhz/msg READ: m0;

Wenn ich die Parameter 

> #define DEBUGDETECT 2
#define DEBUGDECODE 2

einbaue, so schließt er mir sofort die Schnittstelle "closed" und meine TX Lampe bleibt dauerhaft an.

HomeAutoUser, 2017-09-17:
Hier mal noch ein Auszug mit Dataoverflow.
Den Empfänger habe ich abgesteckt und neu angesteckt, da erschien schon anch kurzer Zeit der Fehler.

> 2017.09.17 15:04:16 3: Arduino_Nano_868Mhz/init: disable receiver (XQ)
2017.09.17 15:04:16 5: Arduino_Nano_868Mhz SW: XQ
2017.09.17 15:04:16 4: Arduino_Nano_868Mhz/msg READ: CC1100_PKTCTRL0=50 vs EEPROM PKTCTRL0=50
2017.09.17 15:04:16 4: Arduino_Nano_868Mhz/msg READ: C1100_IOCFG2=13 vs EEPROM IOCFG2=13
2017.09.17 15:04:16 4: Arduino_Nano_868Mhz/msg READ: receiver enabled
2017.09.17 15:04:17 3: Arduino_Nano_868Mhz/init: get version, retry = 0
2017.09.17 15:04:17 5: Arduino_Nano_868Mhz SW: V
2017.09.17 15:04:17 4: Arduino_Nano_868Mhz/msg READ: V 3.3.1-dev_170830_mod_addData SIGNALduino cc1101 - compiled at Sep 17 2017 14:37:19
2017.09.17 15:04:17 5: Arduino_Nano_868Mhz/msg READ: regexp=V\s.*SIGNAL(duino|ESP).* cmd=version msg=V 3.3.1-dev_170830_mod_addData SIGNALduino cc1101 - compiled at Sep 17 2017 14:37:19
2017.09.17 15:04:17 2: Arduino_Nano_868Mhz: initialized. v3.3.1-dev
2017.09.17 15:04:17 5: Arduino_Nano_868Mhz SW: XE
2017.09.17 15:04:17 3: Arduino_Nano_868Mhz/init: enable receiver (XE)
2017.09.17 15:04:21 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1006;LH=931;SL=-518;SH=453;D=6B37D02E1DD6D9B18;C=484;L=65;R=251;
2017.09.17 15:04:21 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1005;LH=921;SL=-525;SH=456;D=A05C3BADBBDB;C=484;L=48;R=251;
2017.09.17 15:04:22 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1022;LH=935;SL=-526;SH=445;D=5BBE8170EEB6DC2C;C=487;L=62;R=251;
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=12 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=59 msglen=61 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-2718;LH=3159;SL=-1224;SH=1683;D=DFB1FF91;C=1463;L=32;R=235;
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=8 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=10 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=7 msglen=10 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=5 msglen=9 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=15 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=17 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=39 msglen=40 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-2720;LH=3171;SL=-1241;SH=1677;D=DFB18;C=1468;L=17;R=235;
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=5 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=4 msglen=4 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=19 msglen=23 bufferMove!
2017.09.17 15:04:28 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=10 bufferMove!
2017.09.17 15:04:30 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=29 bufferMove!
2017.09.17 15:04:30 4: Arduino_Nano_868Mhz/msg READ: start=28 msglen=28 bufferMove!
2017.09.17 15:04:30 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1029;LH=926;SL=-544;SH=425;D=ACD16B3CF828E5571D4F8;C=487;L=81;R=65;
2017.09.17 15:04:30 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1024;LH=928;SL=-539;SH=441;D=A8EB345AEF3E0A3955C644E;C=488;L=91;R=65;
2017.09.17 15:04:30 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1029;LH=926;SL=-535;SH=433;D=AE29974A4183EB8D5470420;C=487;L=90;R=65;
2017.09.17 15:04:38 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1017;LH=934;SL=-536;SH=450;D=AC8860B07558E7D6;C=489;L=63;R=20;
2017.09.17 15:04:39 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1007;LH=938;SL=-521;SH=456;D=45AE8860B07558F6A6;C=486;L=71;R=21;
2017.09.17 15:04:39 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1015;LH=937;SL=-530;SH=452;D=A2D6C430583AAC7783;C=488;L=72;R=20;
2017.09.17 15:04:43 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1003;LH=946;SL=-527;SH=456;D=C7E0A7955B768E;C=488;L=55;R=252;
2017.09.17 15:04:43 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1018;LH=929;SL=-535;SH=436;D=5CFB2E944E07D61AA926000;C=486;L=89;R=252;
2017.09.17 15:04:43 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1010;LH=941;SL=-527;SH=454;D=829E556DF978;C=488;L=45;R=252;
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=14 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=12 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=10 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=10 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=15 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=11 msglen=14 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=7 msglen=9 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=8 msglen=9 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=5 msglen=6 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=11 msglen=14 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=4 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=8 msglen=13 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=10 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=10 msglen=11 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=14 bufferMove!
2017.09.17 15:04:52 4: Arduino_Nano_868Mhz/msg READ: start=6 msglen=7 bufferMove!
[... cut ...]
2017.09.17 15:06:17 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1020;LH=930;SL=-528;SH=443;D=1F829E556DDA38;C=486;L=53;R=252;
2017.09.17 15:06:17 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1017;LH=932;SL=-527;SH=447;D=5AEC7E0A7955B67FE;C=487;L=67;R=252;
2017.09.17 15:06:17 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-1005;LH=953;SL=-528;SH=453;D=2D6E3F053CAADBF2F;C=489;L=68;R=252;

Ich werde sofort nach einem Schälchen Heeßen noch einmal versuchen die Debugausgaben einzufügen welche beim ersten mal sofort dem Empfänger zum "schweigen" brachten.

HomeAutoUser, 2017-09-17:
Hier kommt mal noch ein Auszug, wo man erkennen kann, das der Empfänger kurzzeitig "dich" macht für Befehle, wenn ich das richtig deute.

> 2017.09.17 18:53:55 4: set Arduino_Nano_868Mhz disableMessagetype unsyncedMU CDU
2017.09.17 18:53:56 4: Arduino_Nano_868Mhz/HandleWriteQueue: nothing to send, stopping timer
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=16 bufferMove!
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=13 msglen=14 bufferMove!
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=55 msglen=57 bufferMove!
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=11 bufferMove!
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=14 bufferMove!
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=15 bufferMove!
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=12 msglen=13 bufferMove!
2017.09.17 18:53:58 4: Arduino_Nano_868Mhz/msg READ: start=3 msglen=38 bufferMove!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=60 bufferMove!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: Received answer (start=2 msglen=60 bufferMove!) for config does not match ^MS.*MU.*MC.*
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: start=58 msglen=59 bufferMove!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: Received answer (start=58 msglen=59 bufferMove!) for config does not match ^MS.*MU.*MC.*
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=6 bufferMove!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: Received answer (start=2 msglen=6 bufferMove!) for config does not match ^MS.*MU.*MC.*
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=28 bufferMove!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: Received answer (start=2 msglen=28 bufferMove!) for config does not match ^MS.*MU.*MC.*
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=27 bufferMove!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: Received answer (start=9 msglen=27 bufferMove!) for config does not match ^MS.*MU.*MC.*
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: MS=1;MU=0;MC=0
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/HandleWriteQueue: nothing to send, stopping timer
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=6 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=5 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=6 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=7 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
[... cut ...]
2017.09.17 18:54:14 4: Arduino_Nano_868Mhz/msg READ: start=9 msglen=10 bufferMove!
2017.09.17 18:54:14 4: Arduino_Nano_868Mhz/msg READ: start=121 msglen=122 bufferMove!
2017.09.17 18:54:14 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=12 bufferMove!
2017.09.17 18:54:14 4: Arduino_Nano_868Mhz/msg READ: start=2 msglen=12 bufferMove!
2017.09.17 18:54:14 4: Arduino_Nano_868Mhz/msg READ: start=8 msglen=11 bufferMove!
2017.09.17 18:54:14 4: Arduino_Nano_868Mhz/msg READ: start=6 msglen=6 bufferMove!

Ralf9, 2017-09-17:
Das mit dem BufferMove sieht gut aus.

> Auftreten des add Data overflow! nur bei AKTIVEN MC-Protokoll

Dies passt hier aber nicht. Das "addData overflow" kommt auch bei MU=0;MC=0
```
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: MS=1;MU=0;MC=0
2017.09.17 18:54:00 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow
```

HomeAutoUser, 2017-09-21:
Hallo,

habe ich das richtig verstanden?
> mTrunc=1 bedeuted, daß der buffer voll war und es ein bufferMove gegeben hat. Beim bufferMove müsste normalerweise auch die messageLen verringert werden. Da scheint irgendwas nicht zu passen.

sollte erst auftauchen wenn eine Nachricht empfangen wurde?

Ich habe das bufferMove teilweise auch wenn ich den Empfänger abstecke und nur an ein ein System anstecke, also sofort nach dem CC1101 erkennen.

```
Reading values fom eeprom
CCInit SRES Started
POR Done
CCVersion=20
CCPartnum=0
CC1101 found
Starting timerjob
CC1100_PKTCTRL0=50 vs EEPROM PKTCTRL0=50
C1100_IOCFG2=13 vs EEPROM IOCFG2=13
receiver enabled
start=2 msglen=16 bufferMove!
start=5 msglen=17 bufferMove!
start=15 msglen=19 bufferMove!
start=3 msglen=48 bufferMove!
start=108 msglen=109 bufferMove!
start=2 msglen=10 bufferMove!
start=12 msglen=17 bufferMove!
start=4 msglen=33 bufferMove!
start=4 msglen=163 bufferMove!
start=205 msglen=205 bufferMove!

```

MfG

HomeAutoUser, 2017-09-21:
Soeben habe ich es nochmal mit
```
#define DEBUGDETECT 2
#define DEBUGDECODE 2
```

probiert. Sobald ich an die Schnittstelle gehe 

> ▒Reading values fom eeprom
CCInit SRES Started
POR Done
CCVersion=20
CCPartnum=0
CC1101 found
Starting timerjob
 Clock: 15948, Tol: 150, PattLen: 0 (0), Pulse: 0, 1974, mStart: 0, MCD: ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒

.... und es läuft und läuft ....

Ralf9, 2017-09-21:
> .... und es läuft und läuft ....

Du kannst mal testen ob es besser wird, wenn Du das
#define DEBUGDETECT 2
rausnimmst

Das bufferMove wird auch bei Add pattern verwendet, mir ist aber die funktionsweise nicht klar

```
		// Add pattern
		if (patternLen == maxNumPattern)
		{
			calcHisto();
			if (histo[patternLen] > 2) processMessage();
			for (int16_t i = messageLen - 1; i > 0; --i)
			{
				if (message[i] == pattern_pos) // Finde den letzten Verweis im Array auf den Index der gleich ueberschrieben wird
				{
					i++; // i um eins erhoehen, damit zukuenftigen Berechnungen darauf aufbauen koennen
					bufferMove(i);
					
					break;
				}
			}
}
```

Ralf9, 2017-09-21:
> Ich habe mal noch etwas entlocken können.
> Ob es was brauchbares ist, keine Ahnung.

Es ist wahrscheinlich nur brauchbar, wenn das "addData overflow" dabei vorkommt.

HomeAutoUser, 2017-09-21:
Wenn ich die DEBUGDETECT 2 herausnehme

> Reading values fom eeprom
CCInit SRES Started
POR Done
CCVersion=20
CCPartnum=0
CC1101 found
Starting timerjob
CC1100_PKTCTRL0=50 vs EEPROM PKTCTRL0=50
C1100_IOCFG2=13 vs EEPROM IOCFG2=13
receiver enabled
start=7 msglen=10 bufferMove!
start=7 msglen=16 bufferMove!
 check:mcrst:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst:start=14 msglen=155 bufferMove!
 check:mcrst:start=138 msglen=143 bufferMove!
start=6 msglen=6 bufferMove!
start=2 msglen=18 bufferMove!
start=15 msglen=19 bufferMove!
start=4 msglen=6 bufferMove!
start=6 msglen=9 bufferMove!
start=8 msglen=24 bufferMove!
 check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst:start=18 msglen=142 bufferMove!
 check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst: check:mcrst:val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
 check:mcrst:val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
 check:mcrst:val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
 check:mcrst:val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
[... cut ...]
 check:mcrst:val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
 check:mcrst:val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
 check:mcrst: check:mcrst:val=4 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
 check:mcrst:start=252 msglen=252 bufferMove!
val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=0 addData overflow!!
 check:mcrst: MU found: Mu;▒Ȃ;▒▒;▒▒;▒؉;▒Ч;▒▒;dgUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUP%UUUUUUUUUUUUUUUUUUUuUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU;C5;R0;

Ralf9, 2017-09-21:
ich denke, da kommen wir ohne Sidey nicht weiter.
@sidey79 kannst Du es Dir auch mal anschauen. Hast Du ein Verdacht oder Idee was den addData overflow verursachen könnte?

HomeAutoUser, 2017-09-21:
hier kommt noch eine Variante wo dann der Empfänger den "Empfang einstellt" und nix mehr erkennt.

```
 Clock: 15948, Tol: 6400, PattLen: 8 (3), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
Signal: 46762222222222222222222222222222222222222222222.  [47]
Pattern:  P0: 0*[3073] P1: 0*[-3136] P2: 41*[-32001] P3: 0*[4186] P4: 1*[-349] P5: 0*[5748] P6: 2*[1705] P7: 1*[-687]
 check:mcrst:
  --  chk MC --
mstart:0
0=2,1=6,CLK=6:2,6,nothing to to
Message received:

 Clock: 15948, Tol: 6400, PattLen: 8 (3), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
Signal: 467622222222222222222222222222222222222222222222.  [48]
Pattern:  P0: 0*[3073] P1: 0*[-3136] P2: 44*[-32001] P3: 0*[4186] P4: 1*[-349] P5: 0*[5748] P6: 2*[1705] P7: 1*[-687]
 check:mcrst:
  --  chk MC --
mstart:0
0=2,1=6,CLK=6:2,6,nothing to to
 Clock: 21471, Tol: 6400, PattLen: 8 (3), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
Signal: 467622222222222222222222222222222222222222222222.  [48]
Pattern:  P0: 0*[3073] P1: 0*[-3136] P2: 44*[-32001] P3: 0*[4186] P4: 1*[-349] P5: 0*[5748] P6: 2*[1705] P7: 1*[-687]
 check:mcrst:
  --  chk MC --
mstart:0
0=2,1=6,: 1*:143,CLK=10:0,CLK=10:0,CLK=0,CLK=10:0,CLK=10:0,CLK=10:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=10:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=49:0,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=0:0,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=227:111,CLK=75:111,CLK=75:111,CLK=75:111,CLK=75:111,CLK=75:111,CLK=44:101,CLK=44:101 [... cut ...] CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=44:101,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=48:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=525.▒,CLK=58:254,CLK=58:254,CLK=56:0,CLK=56:0,CLK=56:0,CLK=56:0,CLK=56:0,CLK=56:0,CLK=56:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=53:0,CLK=51:0,CLK=51:0,CLK=51:0,CLK=51:0,CLK=51:0,CLK=51:0,CLK=51:0,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:254,CLK=58:250

```

... schluss für heut

sidey79, 2017-09-22:
Ich habe es noch nicht geschafft, mir das genauer an zu sehen.
Muss mich aktuell leider um etwas anderes kümmern.

sidey79, 2017-09-24:
Ich habe mir das nun einmal ein wenig angesehen:
`check:mcrst:val=5 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 addData overflow!!
`
Ich glaube, es kommt zu solch einer Konstellation nur durch ein vorheriges buffermove. Da scheint dann etwas nicht zu stimmen. Da sollten eigentlich noch Platz für 2 bit sein, die können wir aber nicht mehr adressieren, da unser bytecnt bereits auf 126 steht.

@HomeAutoUser 
Kannst Du mir noch mitteilen, mit welcher Firmware (version / Commit / build Datum) Du diesen Fehler hattest?

Grüße Sidey

HomeAutoUser, 2017-09-24:
@sidey79,
ich werde noch einmal genau schauen, da ich mit 2 FW ständen getestet habe. Ich vermute, es war die letzte dev-r33_cc1101 mir Ralfs Hilfsvariablen. Um allem aus dem Weg zu gehen, werde ich diese nocheinmal unter die Lupe nehmen. Melde mich.

Ralf9, 2017-10-07:
ich habe bei der dev-r33_cc1101 bei addData einige debug Ausgaben ergänzt

HomeAutoUser, 2017-10-07:
@Ralf9 
welche Ausgaben bzw. Zeilen sind wichtig für eventuelle Informationen.
Bei aktiven Debug wird es ja sehr "füllig".

sidey79, 2017-10-07:
@HomeAutoUser 

Ich weiss noch nicht genau, was wir alles brauchen. Ich habe einen Buffer Dump eingebaut, wenn der Fehler auftritt. Wäre gut, wenn Du die aktualisierte Version compilierst und mir hier die Ausgabe schickst.

Buffermove schließe ich als Fehlerquelle mittlerweile eigentlich aus, für mich sieht es eher so aus, als ob processMessage weder eine MC, MS noch MU Ausgabe schafft. mTruncated ist aber gesetzt, das sehen wir.

HomeAutoUser, 2017-10-07:
@sidey79 
Ab sofort läuft Sie.
Ist die Ausgabe Debug 1 unbedingt notwendig? Ich frage aus einfachem Grunde ;-) Ich kann ja förmlich zusehen wie das Logfile wächst bei 2 Empfängern ;-)

sidey79, 2017-10-07:
@HomeAutoUser 

Debug Meldungen sollten mit der Compiler Konfiguration nur kommen, wenn addData fehl schlägt.
Was für Meldungen haut er dir denn in das Log rein? Bei mir kam seit dem commit nicht eine Debug Meldung rein. :(

HomeAutoUser, 2017-10-07:
@sidey79 
Um Missverständnisse auszuschließen, reden wir vom selbigen?
Ich meinte die Debug Ausgabe im FHEM - Empfänger (Debug 1) wo unter anderem auch die Platternlist aufgeführt wird.

Ich würde das System mit Verbose 3 | Debug 0 laufen lassen nach deiner jetzigen Aussage.

sidey79, 2017-10-07:
Ja, in FHEM ist kein Debug notwendig... Verbose 5 wäre aber besser, sonst werden die Seriell ünbergeben Daten nicht angezeigt

HomeAutoUser, 2017-10-07:
@sidey79 

Gesagt getan addData Willkommen, addData erschien schonmal.

> 2017.10.07 22:33:25 4: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, Dropped due to short time or equal msg
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz/msg READ: MS;P0=477;P1=-4054;P2=-2020;P3=-9072;D=03010102010202020102010102010102020101010201020101020202020202020201010102;CP=0;SP=3;R=3;m0;
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 0 dmsg sD16CEB00E000 length 40 RSSI = -72.5
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, test gleich
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, Dropped due to short time or equal msg
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Matched MS Protocol id 68 -> PFR-130
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 68 dmsg sD16CEB00E000 length 40 RSSI = -72.5
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, test gleich
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, Dropped due to short time or equal msg
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz/msg READ: MS;P0=477;P1=-4054;P2=-2020;P3=-9072;D=03010102010202020102010102010102020101010201020101020202020202020201010102;CP=0;SP=3;R=3;
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 0 dmsg sD16CEB00E000 length 40 RSSI = -72.5
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, test gleich
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, Dropped due to short time or equal msg
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Matched MS Protocol id 68 -> PFR-130
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 68 dmsg sD16CEB00E000 length 40 RSSI = -72.5
2017.10.07 22:33:26 5: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, test gleich
2017.10.07 22:33:26 4: Arduino_Nano_433Mhz Dispatch: sD16CEB00E000, Dropped due to short time or equal msg
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: MS;P1=488;P4=-1936;P5=-971;P6=-4000;D=16141415141415151414151515151515151514141514141514141414141515151515151515;CP=1;SP=6;R=216;O;m2;
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz: Found wrong signalpattern, catched 0 bits, aborting demodulation
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz: Matched MS Protocol id 7 -> weatherID7
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 7 dmsg P7#D9806DF00 length 36 RSSI = -94
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz Dispatch: P7#D9806DF00, test ungleich: disabled
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz Dispatch: P7#D9806DF00, -94 dB, dispatch
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz: dispatch P7#D9806DF00
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz SD_WS07_Parse  SD_WS07 (P7#D9806DF00) length: 9
2017.10.07 22:33:29 5: SD_WS07 converted to bits: 11011001 1 000 000001101101 1111 00000000
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz SD_WS07_T decoded protocolid: 7 sensor id=D9, channel=1, temp=10.9, hum=0, bat=ok
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: MS;P1=499;P4=-1932;P5=-971;P6=-4006;D=16141415141415151414151515151515151514141514141514141414141515151515151515;CP=1;SP=6;R=216;m1;
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz: Found wrong signalpattern, catched 0 bits, aborting demodulation
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz: Matched MS Protocol id 7 -> weatherID7
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 7 dmsg P7#D9806DF00 length 36 RSSI = -94
2017.10.07 22:33:29 5: Arduino_Nano_433Mhz Dispatch: P7#D9806DF00, test gleich
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz Dispatch: P7#D9806DF00, Dropped due to short time or equal msg
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: Sync: -4002 -> SyncFact: -0.25, Clock: 15948, Tol: 104, PattLen: 8 (3), Pulse: 524, -948, mStart: 0, MCD: 0, mtrunc: 1
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040504040504040404040505050505050505060404050404050504040505050505050505040405040405040404040405050505050505050604040504040505040405050505050505050404050404050404040404050505050505050506141415141415151414151515151515.  [254]
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 112*[-86] P1: 15*[514] P2: 0*[] P3: 0*[-9124] P4: 51*[-1933] P5: 72*[-948] P6: 4*[-4002] P7: 0*[]
[... cut ...]
2017.10.07 22:33:43 5: Arduino_Nano_868Mhz Dispatch: 810c04xx0909a00117400000A671, -63.5 dB, dispatch
2017.10.07 22:33:43 5: Arduino_Nano_868Mhz: dispatch 810c04xx0909a00117400000A671
2017.10.07 22:33:43 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.10.07 22:33:43 5: Arduino_Nano_868Mhz: Starting demodulation at Position 1
2017.10.07 22:33:43 5: Arduino_Nano_868Mhz: applying postDemodulation
2017.10.07 22:33:43 5: Arduino_Nano_868Mhz: modified value after postDemodulation:

HomeAutoUser, 2017-10-07:
.... und mal noch eine 2. Ausgabe

> 2017.10.07 22:56:13 4: Arduino_Nano_433Mhz: hideki protocol converted to hex: 75EF4675F7BE9557 with 72 bits, messagestart 0
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz Dispatch: P12#75EF4675F7BE9557, test ungleich: disabled
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz Dispatch: P12#75EF4675F7BE9557, -94 dB, dispatch
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: dispatch P12#75EF4675F7BE9557
2017.10.07 22:56:13 4: Hideki_Parse Arduino_Nano_433Mhz incomming P12#75EF4675F7BE9557
2017.10.07 22:56:13 4: Hideki_Parse SensorTyp = 31 decodedString = 7531ca9f19c2bff9
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz decoded Hideki protocol model=Hideki_31, sensor id=31, channel=1, bat=ok, temp=21.9
2017.10.07 22:56:13 5: deviceCode: Hideki_31_1
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 486 RSSI -94 -> tfa 30.3208.0 
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: extracted data 010100011000010001100111011010100011000100001100000101010101101000101011 (bin)
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: protocol does not match return from method: (undef)
2017.10.07 22:56:13 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz/keepalive ok, retry = 0
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1025;LH=919;SL=-546;SH=431;D=AE7B9895AEF3EAA9AC;C=486;L=72;R=215;
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz: Found manchester Protocol id 10 clock 486 RSSI -94.5 -> OSV2o3
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: extracted data 010100011000010001100111011010100101000100001100000101010101011001010011 (bin)
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: protocol does not match return from method: (undef)
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12 clock 486 RSSI -94.5 -> Hideki protocol
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: extracted data 010100011000010001100111011010100101000100001100000101010101011001010011 (bin)
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: protocol does not match return from method: (Start pattern (10101110) not found)
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12.1 clock 486 RSSI -94.5 -> Hideki protocol not invert
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: extracted data 101011100111101110011000100101011010111011110011111010101010100110101100 (bin)
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz: hideki protocol converted to hex: 75EF46B5F7BE556B with 72 bits, messagestart 0
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz Dispatch: P12#75EF46B5F7BE556B, test ungleich: disabled
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz Dispatch: P12#75EF46B5F7BE556B, -94.5 dB, dispatch
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: dispatch P12#75EF46B5F7BE556B
2017.10.07 22:56:13 4: Hideki_Parse Arduino_Nano_433Mhz incomming P12#75EF46B5F7BE556B
2017.10.07 22:56:13 4: Hideki_Parse SensorTyp = 31 decodedString = 7531cadf19c2ffbd
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz decoded Hideki protocol model=Hideki_31, sensor id=31, channel=1, bat=ok, temp=21.9
2017.10.07 22:56:13 5: deviceCode: Hideki_31_1
2017.10.07 22:56:13 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 486 RSSI -94.5 -> tfa 30.3208.0 
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: extracted data 010100011000010001100111011010100101000100001100000101010101011001010011 (bin)
2017.10.07 22:56:13 5: Arduino_Nano_433Mhz: protocol does not match return from method: (undef)
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz/msg READ: MS;P0=468;P1=-9078;P2=-2025;P7=-4056;D=01070702070202020702070702070702020702020707020707020202020202020207020207;CP=0;SP=1;R=59;O;m2;
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1
2017.10.07 22:56:14 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 0 dmsg sD16C9B009000 length 40 RSSI = -44.5
2017.10.07 22:56:14 5: Arduino_Nano_433Mhz Dispatch: sD16C9B009000, test ungleich: disabled
2017.10.07 22:56:14 5: Arduino_Nano_433Mhz Dispatch: sD16C9B009000, -44.5 dB, dispatch
2017.10.07 22:56:14 5: Arduino_Nano_433Mhz: dispatch sD16C9B009000
2017.10.07 22:56:14 4: CUL_TCM97001: W174 original-msg: D16C9B009000 , reversed nibbles: b8639d009000
2017.10.07 22:56:14 4: CUL_TCM97001: W174 nibble 2: 6 , nibble 3: 3
2017.10.07 22:56:14 4: CUL_TCM97001: W174 detected rain gauge message ok
2017.10.07 22:56:14 4: CUL_TCM97001: W174 battery bit: 0
2017.10.07 22:56:14 4: CUL_TCM97001: W174 rain gauge swing count: 217
2017.10.07 22:56:14 4: CUL_TCM97001: W174 rain total: 54.5 l/qm
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz: Matched MS Protocol id 68 -> PFR-130
2017.10.07 22:56:14 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 68 dmsg sD16C9B009000 length 40 RSSI = -44.5
2017.10.07 22:56:14 5: Arduino_Nano_433Mhz Dispatch: sD16C9B009000, test gleich
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz Dispatch: sD16C9B009000, Dropped due to short time or equal msg
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz/msg READ: MS;P0=480;P1=-9075;P2=-2014;P7=-4054;D=01070702070202020702070702070702020702020707020707020202020202020207020207;CP=0;SP=1;R=59;O;m1;
2017.10.07 22:56:14 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1
2017.10.07 22:56:14 5: Arduino_Nano_433Mhz: Starting demodulation at Position 2
[... cut ...]
2017.10.07 22:56:19 4: Arduino_Nano_433Mhz/msg READ: Sync: -4001 -> SyncFact: -0.25, Clock: 15948, Tol: 6400, PattLen: 8 (4), Pulse: -32001, -971, mStart: 0, MCD: 0, mtrunc: 1
2017.10.07 22:56:19 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040504040504040404040505050505050505060404050404050504040505050505050505040405040405040404040405050505050505050604040504040505040405050505050505050404050404050404040404050505050505050506141415141415151414151515151515.  [254]

sidey79, 2017-10-07:
Danke für die Daten. Ich habe jetzt etwas gebastelt und die Daten in den Arduino gebracht :)
Ich verstehe leider noch nicht, wie diese Daten es in dieser Konstellation überhaupt in den message puffer geschafft haben.

Was ich jetzt verifiziert habe ist, dass processMessage die Daten nicht verarbeitet. wäre m_truncated nicht true, dann würde auch ein reset ausgeführt.

Ralf9, 2017-10-07:
@sidey79 
Für mich passt dies nicht richtig zusammen. Wenn state=0 searching ist, dann passt es nicht, daß es eine clock und sync gibt.

```
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ:
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: Sync: -4002 -> SyncFact: -0.25, Clock: 15948, Tol: 104, PattLen: 8 (3), Pulse: 524, -948, mStart: 0, MCD: 0, mtrunc: 1
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040504040504040404040505050505050505060404050404050504040505050505050505040405040405040404040405050505050505050604040504040505040405050505050505050404050404050404040404050505050505050506141415141415151414151515151515. [254]
2017.10.07 22:33:29 4: Arduino_Nano_433Mhz/msg READ: Pattern: P0: 112*[-86] P1: 15*[514] P2: 0*[] P3: 0*[-9124] P4: 51*[-1933] P5: 72*[-948] P6: 4*[-4002] P7: 0*[]

2017.10.07 22:56:17 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.07 22:56:17 4: Arduino_Nano_433Mhz/msg READ:
2017.10.07 22:56:17 4: Arduino_Nano_433Mhz/msg READ: Sync: -4001 -> SyncFact: -0.25, Clock: 15948, Tol: 100, PattLen: 8 (3), Pulse: 504, -971, mStart: 0, MCD: 0, mtrunc: 1
2017.10.07 22:56:17 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040504040504040404040505050505050505060404050404050504040505050505050505040405040405040404040405050505050505050604040504040505040405050505050505050404050404050404040404050505050505050506141415141415151414151515151515. [254]
2017.10.07 22:56:17 4: Arduino_Nano_433Mhz/msg READ: Pattern: P0: 112*[-84] P1: 15*[501] P2: 0*[] P3: 0*[-9108] P4: 51*[-1954] P5: 72*[-971] P6: 4*[-4001] P7: 0*[]

```

sidey79, 2017-10-07:
Lass dich da nicht in die Irre führen. Es gibt weder clock noch sync. Die Variablen sind nur nicht gelöscht. Genauer gesagt, es wird keine clock gefunden und deshalb klappt auch die MU Ausgabe nicht.

Idee: Alles löschen, was bis zu einer möglichen Clock passt.

Ralf9, 2017-10-07:
So langsam kann ich das Problem erahnen,
In die processMessage müsste eine Abfrage rein, etwa so:
```
if (state == searching && messageLen >= maxMsgSize) {
  reset();
  success = true;
```

oder

```
void SignalDetectorClass::processMessage()
{
	if (mcDetected == true || messageLen >= minMessageLen) {
		success = false;
		if (messageLen >= maxMsgSize) {
			m_overflow = true;
			if (state == searching) {
				reset();
  				success = true;
```

sidey79, 2017-10-07:
Das wäre die Holzhammer Methode..... Die Abfrage muss aber am Ende rein und nicht am Anfang :)

sidey79, 2017-10-07:
Also gut, ich habe mir den Datensatz von oben geschnappt und importiert:

`
Sync:  Clock: NULL, Tol: 150, PattLen: 8 (8), Pulse: 514, -948, mStart: 0, MCD: 0, mtrunc: 1
Signal: 06040405040405050404050505050505050504040504040504040404040505050505050505060404050404050504040505050505050505040405040405040404040405050505050505050604040504040505040405050505050505050404050404050404040404050505050505050506141415141415151414151515151515.  [254]
Pattern:  P0: 112*[-86] P1: 15*[514] P2: 0*[] P3: 0*[] P4: 51*[-1933] P5: 72*[-948] P6: 4*[-4002] P7: 0*[]
`
nach processMessage() sieht dieser Datensatz dann so aus:
`
Sync:  Clock: NULL, Tol: 102, PattLen: 8 (8), Pulse: 514, -948, mStart: 0, MCD: 0, mtrunc: 1
Signal: 61414151414151514141515151515151.  [32]
Pattern:  P0: 112*[-86] P1: 15*[514] P2: 0*[] P3: 0*[] P4: 51*[-1933] P5: 72*[-948] P6: 4*[-4002] P7: 0*[]
`
Die Statistik wird nicht neu berechnet, davon mal nicht beirren lassen, Wichtig ist, dass ich das Signal gekürzt habe und zwar bis zu dem Punkt, wo überhaupt erst was valides beginnt.
Wie es diese hintereinander folgenden negativen Pulse in den Puffer geschafft haben, ist mir noch immer unklar.

Ralf9, 2017-10-08:
In Zeile 439 bis 448 wird die Bedingung (success == false && m_truncated == true) auch erfüllt, ist dies so ok?
"Start found, but no end. We remove everything bevore start and hope to find the end later"

sidey79, 2017-10-08:
kannst Du das Mal verlinken was Du meinst? ich stehe irgendwie auf dem Schlauch und finde es nicht.

Ralf9, 2017-10-08:
ich meine das hier in der signaldecoder.cpp
```

			else if (m_endfound == false && mstart > 0 && mend + 1 >= maxMsgSize) // Start found, but no end. We remove everything bevore start and hope to find the end later
			{
				//MSG_PRINT("copy");
#ifdef DEBUGDECODE
				DBG_PRINT(" move msg ");;
#endif
				bufferMove(mstart);
				mstart = 0;
				//m_truncated = true;  // Flag that we truncated the message array and want to receiver some more data
} 
```

HomeAutoUser, 2017-10-08:
Das ganze liest sich nach einem "Kettenproblem".
Ich habe nun seit einer Stunde den Test begonnen mit der neuen FW. Es schaut erstmal gut aus vorerst aber ich möchte nich den Tag vor dem Abend loben.

HomeAutoUser, 2017-10-08:
@sidey79 leider gibt es noch keine Erfolge zu vermelden. Der gleiche Zustand trat wieder auf.
Hier erneut 2 Auszüge.

>2017.10.08 12:23:19 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 487 RSSI -46.5 -> tfa 30.3208.0 
2017.10.08 12:23:25 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-32001;P1=530;P2=-9030;P3=-4163;P4=-2093;D=1213141314131414131314141414141414141313131314131413141313141413131313141413;CP=1;SP=2;R=232;O;m2;
2017.10.08 12:23:25 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1
2017.10.08 12:23:25 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 0 dmsg sA9807AB3C800 length 40 RSSI = -86
2017.10.08 12:23:25 4: CUL_TCM97001 using longid: SD_WS,SD_WS07_TH,14_CUL_TCM97001 model: GT_WT_02
2017.10.08 12:23:25 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 12:23:25 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 12:23:25 4: Arduino_Nano_433Mhz/msg READ: Sync: -9043 -> SyncFact: -0.57, Clock: NULL, Tol: 95, PattLen: 8 (2), Pulse: 476, -2010, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Signal: 02030403040304040303040404040404040403030303040304030403030404030303030404030204020304030403040403030404040404040404030303030403040304030304040303030304040302040203040304056703030403040404030403030403030404030303040303030304040404040404040304040302030304.  [254]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 126*[-6] P1: 0*[476] P2: 6*[-9043] P3: 57*[-4058] P4: 62*[-2010] P5: 1*[-996] P6: 1*[] P7: 1*[-13740]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: val=4 mstart=0 mend=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Sync: -9043 -> SyncFact: -0.57, Clock: NULL, Tol: 409, PattLen: 8 (2), Pulse: -2048, -2029, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Signal: 02030403040304040303040404040404040403030303040304030403030404030303030404030204020304030403040403030404040404040404030303030403040304030304040303030304040302040203040304056703030403040404030403030403030404030303040303030304040404040404040304040302030304.  [254]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 126*[-6] P1: 0*[476] P2: 6*[-9043] P3: 57*[-4058] P4: 62*[-2029] P5: 1*[-996] P6: 1*[] P7: 1*[-13740]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Sync: -9043 -> SyncFact: -0.57, Clock: NULL, Tol: 91, PattLen: 8 (2), Pulse: 456, -2029, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Signal: 02030403040304040303040404040404040403030303040304030403030404030303030404030204020304030403040403030404040404040404030303030403040304030304040303030304040302040203040304056703030403040404030403030403030404030303040303030304040404040404040304040302030304.  [254]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 126*[-6] P1: 0*[466] P2: 6*[-9043] P3: 57*[-4058] P4: 62*[-2029] P5: 1*[-996] P6: 1*[] P7: 1*[-13740]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: val=4 mstart=0 mend=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Sync: -9043 -> SyncFact: -0.57, Clock: NULL, Tol: 404, PattLen: 8 (2), Pulse: -2024, -2026, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Signal: 02030403040304040303040404040404040403030303040304030403030404030303030404030204020304030403040403030404040404040404030303030403040304030304040303030304040302040203040304056703030403040404030403030403030404030303040303030304040404040404040304040302030304.  [254]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 126*[-6] P1: 0*[466] P2: 6*[-9043] P3: 57*[-4058] P4: 62*[-2026] P5: 1*[-996] P6: 1*[] P7: 1*[-13740]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Sync: -9043 -> SyncFact: -0.57, Clock: NULL, Tol: 96, PattLen: 8 (2), Pulse: 480, -2026, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Signal: 02030403040304040303040404040404040403030303040304030403030404030303030404030204020304030403040403030404040404040404030303030403040304030304040303030304040302040203040304056703030403040404030403030403030404030303040303030304040404040404040304040302030304.  [254]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 126*[-6] P1: 0*[473] P2: 6*[-9043] P3: 57*[-4058] P4: 62*[-2026] P5: 1*[-996] P6: 1*[] P7: 1*[-13740]
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: val=4 mstart=0 mend=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 12:23:26 4: Arduino_Nano_433Mhz/msg READ: Sync: -9043 -> SyncFact: -0.57, Clock: NULL, Tol: 404, PattLen: 8 (2), Pulse: -2024, -2025, mStart: 0, MCD: 0, mtrunc: 1
[... cut ...]
2017.10.08 12:47:35 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.10.08 12:47:35 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.10.08 12:47:35 3: OREGON: Unknown device THR128_04_1, please define it
2017.10.08 12:47:48 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-32001;P1=365;P2=-5128;P3=136;P4=-367;P5=851;P7=-854;D=01234545454545417175454541754545454171717175417545454541754175454175417545417175454171717541754175417541717175454171;CP=1;R=3;

sidey79, 2017-10-08:
@HomeAutoUser 
So nach ca. 5 Stunden laufzeit hatte ich das Problem jetzt heute Nacht auch und dann ein paar mal im Verlauf des Tages.
Nachstellen ließ sich genau dieser Fehler nun nicht mehr :(

Da hier Speicherbereiche überschrieben werden und das immer murks ist, habe ich noch was korrigiert, was mir entfallen war.

Am besten noch mal compilieren und beobachten. Die Daten müssen nach der Aufnahme im Puffer so verändert worden sein.

HomeAutoUser, 2017-10-08:
@sidey79 
ich habe schon mitbekommen das du aktiv warst und habe die noch "warme" Variante schon in die Empfänger übertragen. Wir werden das Problem schon finden !

HomeAutoUser, 2017-10-08:
**Wieder nix....**
Ist es Zufall, das es bei einem bestimmten Protokoll vorher auftritt oder sehe ich das nur weil es schon spät ist ;-) ?

>2017.10.08 21:28:02 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 487 RSSI -67.5 -> tfa 30.3208.0 
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz/msg READ: MS;P1=496;P3=-1939;P4=-965;P5=-3983;D=15131314131314141313141414141414141413141313131413131313131414141414141414;CP=1;SP=5;R=221;O;m2;
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz: Matched MS Protocol id 7 -> weatherID7
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 7 dmsg P7#D9805DF00 length 36 RSSI = -91.5
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz SD_WS07_Parse  SD_WS07 (P7#D9805DF00) length: 9
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz SD_WS07_T decoded protocolid: 7 sensor id=D9, channel=1, temp=9.3, hum=0, bat=ok
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz/msg READ: MS;P1=499;P3=-1932;P4=-959;P5=-3990;D=15131314131314141313141414141414141413141313131413131313131414141414141414;CP=1;SP=5;R=221;O;m1;
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz: Matched MS Protocol id 7 -> weatherID7
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 7 dmsg P7#D9805DF00 length 36 RSSI = -91.5
2017.10.08 21:28:04 4: Arduino_Nano_433Mhz Dispatch: P7#D9805DF00, Dropped due to short time or equal msg
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: MS;P1=516;P3=-1937;P4=-952;P5=-3989;D=15131314131314141313141414141414141413141313131413131313131414141414141414;CP=1;SP=5;R=221;m0;
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz: Matched MS Protocol id 7 -> weatherID7
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 7 dmsg P7#D9805DF00 length 36 RSSI = -91.5
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz Dispatch: P7#D9805DF00, Dropped due to short time or equal msg
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Sync: -3984 -> SyncFact: -0.25, Clock: NULL, Tol: 101, PattLen: 8 (2), Pulse: 508, -965, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Signal: 05030304030304040303040404040404040403040303030403030303030404040404040404050303040303040403030404040404040404030403030304030303030304040404040404040503030403030404030304040404040404040304030303040303030303040404040404040405030304030304040303040404040404.  [254]
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[508] P2: 0*[-488] P3: 51*[-1933] P4: 72*[-965] P5: 4*[-3984] P6: 0*[] P7: 0*[]
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: val=4 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Sync: -3984 -> SyncFact: -0.25, Clock: NULL, Tol: 216, PattLen: 8 (2), Pulse: -1080, -1022, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Signal: 05030304030304040303040404040404040403040303030403030303030404040404040404050303040303040403030404040404040404030403030304030303030304040404040404040503030403030404030304040404040404040304030303040303030303040404040404040405030304030304040303040404040404.  [254]
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[508] P2: 0*[-488] P3: 51*[-1933] P4: 72*[-1022] P5: 4*[-3984] P6: 0*[] P7: 0*[]
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: val=2 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Sync: -3984 -> SyncFact: -0.25, Clock: NULL, Tol: 349, PattLen: 8 (3), Pulse: 1748, -1022, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Signal: 05030304030304040303040404040404040403040303030403030303030404040404040404050303040303040403030404040404040404030403030304030303030304040404040404040503030403030404030304040404040404040304030303040303030303040404040404040405030304030304040303040404040404.  [254]
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[508] P2: 0*[1748] P3: 51*[-1933] P4: 72*[-1022] P5: 4*[-3984] P6: 0*[] P7: 0*[]
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: val=4 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Sync: -3984 -> SyncFact: -0.25, Clock: NULL, Tol: 249, PattLen: 8 (3), Pulse: -1248, -1135, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 21:28:05 4: Arduino_Nano_433Mhz/msg READ: Signal: 05030304030304040303040404040404040403040303030403030303030404040404040404050303040303040403030404040404040404030403030304030303030304040404040404040503030403030404030304040404040404040304030303040303030303040404040404040405030304030304040303040404040404.  [254]
[... cut ...]
2017.10.08 22:38:24 4: Arduino_Nano_433Mhz/msg READ: Sync: -3986 -> SyncFact: -0.25, Clock: NULL, Tol: 6400, PattLen: 8 (3), Pulse: -32001, -957, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 22:38:24 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-29328;P1=380;P2=-414;P3=595;P4=-594;D=0121212121212121212121212343412121234343412123434341212341212123434343434121234123412121212123412123434343434121234121;CP=1;R=13;
2017.10.08 22:38:24 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate

HomeAutoUser, 2017-10-09:
@sidey79 
einen Zusammenhang mit dem vermuteten Protocoll 7 hat es nichts zu tun, dies war ein Holzweg. Ich blockte dies und der Fehler kam leider erneut :-(

>2017.10.08 23:56:32 4: Arduino_Nano_433Mhz/msg READ: MS;P0=495;P1=-976;P2=-1946;P3=-3895;D=03020101010101010101010101010101010102010102010202020202020101010102010102;CP=0;SP=3;R=235;m0;
2017.10.08 23:56:32 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.08 23:56:33 4: Arduino_Nano_433Mhz/msg READ: val=2 mstart=0 mend=73 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 state=2 addData overflow!!
2017.10.08 23:56:33 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.08 23:56:33 4: Arduino_Nano_433Mhz/msg READ: Sync: -4048 -> SyncFact: -8.47, Clock: 478, Tol: 406, PattLen: 7 (7), Pulse: -2032, 478, mStart: 0, MCD: 0, mtrunc: 1
2017.10.08 23:56:33 4: Arduino_Nano_433Mhz/msg READ: Signal: 030201010101010101010101010101010101020101020102020202020201010101020101024050503060303020302020203020303020303020203030302030303030202020202020202030202030603030203020202030203030203030202030303020303030302020202020202020302020306030302030202020302030.  [252]
2017.10.08 23:56:33 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 36*[478] P1: 24*[-976] P2: 10*[-2028] P3: 1*[-4048] P4: 0*[132] P5: 0*[-32001] P6: 0*[-9074]
2017.10.08 23:56:33 4: Arduino_Nano_433Mhz/msg READ: MS;P0=478;P2=-2028;P3=-4048;P6=-9074;D=06030302030202020302030302030302020303030203030303020202020202020203020203;CP=0;SP=6;R=235;
2017.10.08 23:56:33 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1

>2017.10.09 00:11:22 4: Arduino_Nano_433Mhz: Fingerprint for MU Protocol id 9 -> CTW 600 matches, trying to demodulate
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz/msg READ: MS;P1=512;P3=-1933;P4=-950;P5=-3998;D=15131314131314141313141414141414141413141313141414131313131414141414141414;CP=1;SP=5;R=223;O;m2;
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz/msg READ: MS;P1=506;P3=-1918;P4=-953;P5=-3982;D=15131314131314141313141414141414141413141313141414131313131414141414141414;CP=1;SP=5;R=223;O;m1;
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz/msg READ: MS;P1=526;P3=-1931;P4=-945;P5=-3977;D=15131314131314141313141414141414141413141313141414131313131414141414141414;CP=1;SP=5;R=223;m0;
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz/msg READ: val=4 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 00:11:28 4: Arduino_Nano_433Mhz/msg READ: Sync: -3996 -> SyncFact: -0.25, Clock: NULL, Tol: 189, PattLen: 8 (3), Pulse: -948, -7, mStart: 0, MCD: 0, mtrunc: 1
2017.10.09 00:11:29 4: Arduino_Nano_433Mhz/msg READ: Signal: 51313141313141413131414141414141414131413131414141313131314141414141414141513131413131414131314141414141414141314131314141413131313141414141414141415131314131314141313141414141414141413141313141414131313131414141414141414151313141313141413131414141414141.  [254]
2017.10.09 00:11:29 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 0*[-61] P1: 127*[-7] P2: 0*[] P3: 45*[-1944] P4: 78*[-947] P5: 4*[-3996] P6: 0*[] P7: 0*[]
2017.10.09 00:11:29 4: Arduino_Nano_433Mhz/msg READ: MU;P1=-7;P3=520;P4=-1943;P5=-952;D=134353534345535353434343435353535353535353;CP=3;R=3;

>2017.10.09 00:12:43 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12.1 clock 488 RSSI -92 -> Hideki protocol not invert
2017.10.09 00:12:43 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 488 RSSI -92 -> tfa 30.3208.0 
2017.10.09 00:12:49 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-20208;P1=391;P2=-402;P3=604;P4=-615;P5=-8956;D=01212121212121212121212123412121234123434341212341212121212123412121212121212121234123412123434121212121212121212121212121212341212341215121212121212121212121212341212123412343434121234121212121212341212121212121212123412341212343412121212121212121212121;CP=1;R=28;O;
2017.10.09 00:12:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.10.09 00:12:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.10.09 00:12:49 4: Arduino_Nano_868Mhz: FHT80 - roomthermostat protolength 54
2017.10.09 00:12:49 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 73 dmsg 810c04xx0909a00117400000A600 length 48 RSSI = -60
2017.10.09 00:12:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.10.09 00:12:58 4: Arduino_Nano_433Mhz/msg READ: val=0 mstart=3 mend=107 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.09 00:12:58 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 00:12:58 4: Arduino_Nano_433Mhz/msg READ: Sync:  Clock: NULL, Tol: 6400, PattLen: 8 (0), Pulse: -32001, -32001, mStart: 3, MCD: 0, mtrunc: 1
2017.10.09 00:12:58 4: Arduino_Nano_433Mhz/msg READ: Signal: 50000000000000000000000000000000000000000000000000000000000000000000000000000000000070000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.  [254]
2017.10.09 00:12:58 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 252*[-32001] P1: 0*[958] P2: 0*[-996] P3: 0*[-502] P4: 0*[477] P5: 1*[92] P6: 0*[-280] P7: 1*[-4104]
[... cut ...]
2017.10.09 00:13:00 4: Arduino_Nano_433Mhz/msg READ: Sync:  Clock: NULL, Tol: 102, PattLen: 8 (0), Pulse: -512, -32001, mStart: 3, MCD: 0, mtrunc: 1
2017.10.09 00:13:02 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-7228;P1=854;P2=-367;P3=362;P4=-866;D=01212121212121212121234341212123434121212341234121234123412123412341212343412121234121234123434123412341234121234341212343;CP=3;R=3;
2017.10.09 00:13:05 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.10.09 00:13:07 4: Arduino_Nano_433Mhz/keepalive ok, retry = 0

Ralf9, 2017-10-09:
> 2017.10.08 23:56:32 4: READ: �MS;P0=495;P1=-976;P2=-1946;P3=-3895;D=03020101010101010101010101010101010102010102010202020202020101010102010102;CP=0;SP=3;R=235;m0;�
> 2017.10.08 23:56:32 4: Matched MS Protocol id 1 -> ConradRSL
> 2017.10.08 23:56:33 4: READ: val=2 mstart=0 mend=73 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 state=2 addData overflow!!
> 2017.10.08 23:56:33 4: READ: Sync: -4048 -> SyncFact: -8.47, Clock: 478, Tol: 406, PattLen: 7 (7), Pulse: -2032, 478, mStart: 0, MCD: 0, mtrunc: 1
> 2017.10.08 23:56:33 4: READ: Signal: 030201010101010101010101010101010101020101020102020202020201010101020101024050503060303020302020203020303020303020203030302030303030202020202020202030202030603030203020202030203030203030202030303020303030302020202020202020302020306030302030202020302030. [252]
> 2017.10.08 23:56:33 4: READ: Pattern: P0: 36*[478] P1: 24*[-976] P2: 10*[-2028] P3: 1*[-4048] P4: 0*[132] P5: 0*[-32001] P6: 0*[-9074]

Dies hier fällt aus der Reihe, dort ist state=2 (syncfound).
Hier ist die Wiederholung von der Nachricht davor (MS Protocol id 1 -> ConradRSL) enthalten.
Mir ist nicht klar warum diese MS-Nachricht nicht ausgegeben wurde.

Hier wird auch ein bufferMove(i) verwendet und m_truncated = true gesetzt. Kann dieses gesetztes m_truncated irgendwelche Nebenwirkungen haben?
https://github.com/RFD-FHEM/SIGNALDuino/blob/bc29fd6abb7bfd52311dd50c740ab11bb504febb/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L149

Ralf9, 2017-10-09:
Ich habe hier einige debuginfos ergänzt 
und bei  doDetect "bufferMove" durch "message.moveLeft" ersetzt, damit sollte es keine addData overflow mit state=2 mehr geben
https://github.com/RFD-FHEM/SIGNALDuino/pull/66/commits/33d277add3191701c820969ac23dfc568d9b06eb

sidey79, 2017-10-09:
@Ralf9 
Was soll das bringen, bufferMove durch message.moveLeft zu ersetzen?
Wenn der Wertebereich valide ist, wird doch ein moveLeft durch bufferMove ausgeführt.

Ich habe allerdings noch einen echten Fehler gefunden:

Der Array `histo` wird mit histo[maxNumPattern] angekegt. Also gibt es histo[0..7]!

Und hier passiert der Fehler, wir greifen auf histo[8] zu...
https://github.com/RFD-FHEM/SIGNALDuino/blob/bc29fd6abb7bfd52311dd50c740ab11bb504febb/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L143

Ralf9, 2017-10-09:
> Was soll das bringen, bufferMove durch message.moveLeft zu ersetzen?
> Wenn der Wertebereich valide ist, wird doch ein moveLeft durch bufferMove ausgeführt.

Ich möchte damit verhindern, daß beim doDetect "bufferMove" auch das "m_truncated = true" gesetzt wird

sidey79, 2017-10-09:
Naja, die Idee von m_truncated ist ja, dass es gesetzt wird, nachdem der Puffer verschoben wurde.
Bin ich jetzt noch nicht so begeistert, wenn man sich nicht darauf verlassen kann, wenn diese Flags gesetzt werden

Ralf9, 2017-10-09:
Das m_truncated wird benötigt, daß nach einem bufferMove in der processMessage am Ende kein reset() durchgeführt wird

Ralf9, 2017-10-09:
Ich hatte die Vermutung, daß durch das "m_truncated = true im  doDetect "bufferMove" im processMessage ein voller message Buffer nicht gelöscht wird, oder habe ich da was übersehen?
Kann der histo[8] Fehler der Grund für diesen state=2 addData overflow sein?
> 2017.10.08 23:56:33 4: READ: val=2 mstart=0 mend=73 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 state=2 addData overflow!!

HomeAutoUser, 2017-10-09:
Guten Abend,

hier kommt wieder etwas Lektüre nachdem ich die aktuelle von @sidey79 bereitgestellte Version in den Empfänger geschoben habe.

>2017.10.09 23:56:19 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 488 RSSI -45.5 -> tfa 30.3208.0 
2017.10.09 23:56:20 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.10.09 23:56:20 4: Arduino_Nano_433Mhz/keepalive ok, retry = 0
2017.10.09 23:56:24 4: Arduino_Nano_433Mhz/msg READ: MS;P1=511;P4=-1931;P5=-947;P6=-3992;D=16141415141415151414151515151515151514141414151514141414141515151515151515;CP=1;SP=6;R=221;O;m2;
2017.10.09 23:56:24 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 107796 mb f a t proc 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Sync: -3986 -> SyncFact: -0.25, Clock: NULL, Tol: 215, PattLen: 8 (1), Pulse: 380, -1011, mStart: 0, MCD: 0, mtrunc: 1
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040404050504040404040505050505050505060404050404050504040505050505050505040404040505040404040405050505050505050604040504040505040405050505050505050404040405050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-4] P1: 0*[] P2: 0*[-472] P3: 0*[] P4: 51*[-1933] P5: 72*[-1011] P6: 4*[-3986] P7: 0*[]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: val=1 mstart=0 mend=150 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Sync: -3986 -> SyncFact: -0.25, Clock: NULL, Tol: 76, PattLen: 8 (2), Pulse: 380, -1011, mStart: 0, MCD: 0, mtrunc: 1
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040404050504040404040505050505050505060404050404050504040505050505050505040404040505040404040405050505050505050604040504040505040405050505050505050404040405050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-4] P1: 0*[380] P2: 0*[-472] P3: 0*[] P4: 51*[-1933] P5: 72*[-1011] P6: 4*[-3986] P7: 0*[]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 107980 mb f a t proc 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Sync: -3986 -> SyncFact: -0.25, Clock: NULL, Tol: 76, PattLen: 8 (2), Pulse: -964, -1011, mStart: 0, MCD: 0, mtrunc: 1
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040404050504040404040505050505050505060404050404050504040505050505050505040404040505040404040405050505050505050604040504040505040405050505050505050404040405050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-4] P1: 0*[380] P2: 0*[-472] P3: 0*[] P4: 51*[-1933] P5: 72*[-1011] P6: 4*[-3986] P7: 0*[]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: val=5 mstart=0 mend=150 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Sync: -3986 -> SyncFact: -0.25, Clock: NULL, Tol: 192, PattLen: 8 (2), Pulse: -964, -987, mStart: 0, MCD: 0, mtrunc: 1
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040404050504040404040505050505050505060404050404050504040505050505050505040404040505040404040405050505050505050604040504040505040405050505050505050404040405050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-4] P1: 0*[380] P2: 0*[-472] P3: 0*[] P4: 51*[-1933] P5: 72*[-987] P6: 4*[-3986] P7: 0*[]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 108172 mb f a t proc 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Sync: -3986 -> SyncFact: -0.25, Clock: NULL, Tol: 192, PattLen: 8 (2), Pulse: 508, -987, mStart: 0, MCD: 0, mtrunc: 1
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040404050504040404040505050505050505060404050404050504040505050505050505040404040505040404040405050505050505050604040504040505040405050505050505050404040405050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-4] P1: 0*[380] P2: 0*[-472] P3: 0*[] P4: 51*[-1933] P5: 72*[-987] P6: 4*[-3986] P7: 0*[]
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: val=2 mstart=0 mend=150 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Sync: -3986 -> SyncFact: -0.25, Clock: NULL, Tol: 101, PattLen: 8 (3), Pulse: 508, -987, mStart: 0, MCD: 0, mtrunc: 1
2017.10.09 23:56:25 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050504040404050504040404040505050505050505060404050404050504040505050505050505040404040505040404040405050505050505050604040504040505040405050505050505050404040405050404040404050505050505050506040405040405050404050505050505.  [254]
[... cut ...]
2017.10.09 23:56:49 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-32001;P1=851;P2=-367;P3=361;P4=-862;D=01212121212121212121234341212123434121212341212341234123412123412341212341212121234343412123434123412341234121234341212343;CP=3;R=38;
2017.10.09 23:56:54 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-32001;P1=393;P2=-401;P3=612;P4=-617;P5=-8924;D=01212121212121212121212123412121234123434341212341212121212123412121212121212121234123412123434121212121212121212121212121212341212341215121212121212121212121212341212123412343434121234121212121212341212121212121212123412341212343412121212121212121212121;CP=1;R=26;O;

sidey79, 2017-10-10:
meine Ergänzenden Ausgaben haben leider nichts gebracht. Was bedeutet, dass dort nicht das Problem liegt.

Irgendwo passiert aber was mit dem pattern Puffer. 
Scheint echt aufwändig zu werden. Wenn ich den Fehler doch nur simulieren könnte

HomeAutoUser, 2017-10-10:
Hier mal noch eine Ausgabe vom erstmaligen auftreten beim 868Mhz Empfänger.

>2017.10.10 15:59:28 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-26016;P1=397;P2=-395;P3=619;P4=-601;P5=-8984;D=01212121212121212121212123412121234123434123412121234123434341212121212121212121234123412123434121212121212121212121234341234343434343415121212121212121212121212341212123412343412341212123412343434121212121212121212123412341212343412121212121212121212123;CP=1;R=11;O;
2017.10.10 15:59:28 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.10.10 15:59:28 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.10.10 15:59:28 4: Arduino_Nano_868Mhz: FHT80 - roomthermostat protolength 54
2017.10.10 15:59:28 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 73 dmsg 810c04xx0909a00116170000A600 length 48 RSSI = -68.5
2017.10.10 15:59:28 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.10.10 15:59:43 4: Arduino_Nano_433Mhz/keepalive ok, retry = 0
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 57897962 mb f a t proc 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 6400, PattLen: 8 (1), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Signal: 54666666666666666666666666706666666666666666666666666666666666666666670666666666666666666666666606666666666666666666666666666666666666666666666666666666666666666660666666666666666666666666666666666666666666666666666666666666666666066666606666666666666666.  [254]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 6*[7859] P1: 0*[-364] P2: 0*[160] P3: 0*[-116] P4: 1*[113] P5: 1*[-156] P6: 244*[-32001] P7: 2*[-16220]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: val=6 mstart=0 mend=0 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 6400, PattLen: 8 (1), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Signal: 54666666666666666666666666706666666666666666666666666666666666666666670666666666666666666666666606666666666666666666666666666666666666666666666666666666666666666660666666666666666666666666666666666666666666666666666666666666666666066666606666666666666666.  [254]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 6*[7859] P1: 0*[-364] P2: 0*[160] P3: 0*[-116] P4: 1*[113] P5: 1*[-156] P6: 244*[-32001] P7: 2*[-16220]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 57898142 mb f a t proc 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 6400, PattLen: 8 (1), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Signal: 54666666666666666666666666706666666666666666666666666666666666666666670666666666666666666666666606666666666666666666666666666666666666666666666666666666666666666660666666666666666666666666666666666666666666666666666666666666666666066666606666666666666666.  [254]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 6*[7859] P1: 0*[-364] P2: 0*[160] P3: 0*[-116] P4: 1*[113] P5: 1*[-156] P6: 244*[-32001] P7: 2*[-16220]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: val=6 mstart=0 mend=0 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 6400, PattLen: 8 (1), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Signal: 54666666666666666666666666706666666666666666666666666666666666666666670666666666666666666666666606666666666666666666666666666666666666666666666666666666666666666660666666666666666666666666666666666666666666666666666666666666666666066666606666666666666666.  [254]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 6*[7859] P1: 0*[-364] P2: 0*[160] P3: 0*[-116] P4: 1*[113] P5: 1*[-156] P6: 244*[-32001] P7: 2*[-16220]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 57898330 mb f a t proc 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 6400, PattLen: 8 (1), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Signal: 54666666666666666666666666706666666666666666666666666666666666666666670666666666666666666666666606666666666666666666666666666666666666666666666666666666666666666660666666666666666666666666666666666666666666666666666666666666666666066666606666666666666666.  [254]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 6*[7859] P1: 0*[-364] P2: 0*[160] P3: 0*[-116] P4: 1*[113] P5: 1*[-156] P6: 244*[-32001] P7: 2*[-16220]
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: val=6 mstart=0 mend=0 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.10 15:59:47 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 6400, PattLen: 8 (1), Pulse: -32001, -32001, mStart: 0, MCD: 0, mtrunc: 1
[... cut ...]
2017.10.10 15:59:49 4: Arduino_Nano_868Mhz/msg READ: Signal: 54666666666666666666666666706666666666666666666666666666666666666666670666666666666666666666666606666666666666666666666666666666666666666666666666666666666666666660666666666666666666666666666666666666666666666666666666666666666666066666606666666666666666.  [254]
2017.10.10 16:00:24 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.10.10 16:00:43 4: Arduino_Nano_433Mhz/KeepAlive not ok, retry = 1 -> get ping
2017.10.10 16:00:44 4: Arduino_Nano_433Mhz/msg READ: OK

sidey79, 2017-10-10:
Ich denke ich bin dem Problem ein kleines bisschen Näher gekommen.
Das Problem liegt vermutlich doch in der Bitstore Klasse...

Es sind 252 Bits im Message Puffer. Allerdings sind die Variablen in der Bitstore Klasse so gesetzt, dass die Bitsore Klasse keine neuen Werte mehr annimmt. So auf den 1. Blick stimmen alle Werte bis auf bytecnt.
Bytecnt müsste 125 sein, das wäre dann genau 126 bit mit je zwei Werten = 252 Einträge im Puffer:

`
OOB
bcnt=7val=7 mstart=0 mend=0 msglen=252 bytecnt=126 valcnt=252 mTrunc=1 state=3 addData overflow!!

Sync:  Clock: NULL, Tol: 118, PattLen: 8 (5), Pulse: 592, -173, mStart: 0, MCD: 0, mtrunc: 1
Signal: 777007071707070707002525257025702570257025252570237071707070707070707025252570457025702570252525702371707070707070707070252525702570257025702525257023707070707070707070702525257025702570257045252570237070707070707070707025252570257025702570254525704371.  [252]
Pattern:  P0: 23*[-224] P1: 2*[-173] P2: 13*[180] P3: 1*[-5816] P4: 0*[141] P5: 12*[-585] P6: 0*[100] P7: 25*[580]
`

HomeAutoUser, 2017-10-10:
@sidey79,

dann kannst du hier schauen ob deine Erkenntnisse sich bestätigen. Deine neue FW wurde reingescchoben und nach wenigen Minuten das ...

>2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: MS;P1=514;P4=-1930;P5=-949;P6=-4006;D=16141415141415151414151515151515151415151514141514141414141515151515151515;CP=1;SP=6;R=219;O;m2;
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: MS;P1=502;P4=-1933;P5=-964;P6=-3996;D=16141415141415151414151515151515151415151514141514141414141515151515151515;CP=1;SP=6;R=219;O;m1;
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz: Matched MS Protocol id 1 -> ConradRSL
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: 76766 mb f a t proc 
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Sync: -3996 -> SyncFact: -0.25, Clock: NULL, Tol: 188, PattLen: 8 (1), Pulse: 528, -957, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050405050504040504040404040505050505050505060404050404050504040505050505050504050505040405040404040405050505050505050604040504040505040405050505050505040505050404050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[] P2: 0*[-480] P3: 0*[-9096] P4: 48*[-1932] P5: 75*[-957] P6: 4*[-3996] P7: 0*[]
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: OOB
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: bcnt=7val=1 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Sync: -3996 -> SyncFact: -0.25, Clock: NULL, Tol: 105, PattLen: 8 (2), Pulse: 528, -957, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050405050504040504040404040505050505050505060404050404050504040505050505050504050505040405040404040405050505050505050604040504040505040405050505050505040505050404050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[528] P2: 0*[-480] P3: 0*[-9096] P4: 48*[-1932] P5: 75*[-957] P6: 4*[-3996] P7: 0*[]
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: 76952 mb f a t proc 
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Sync: -3996 -> SyncFact: -0.25, Clock: NULL, Tol: 105, PattLen: 8 (2), Pulse: -940, -957, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 21:56:51 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050405050504040504040404040505050505050505060404050404050504040505050505050504050505040405040404040405050505050505050604040504040505040405050505050505040505050404050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[528] P2: 0*[-480] P3: 0*[-9096] P4: 48*[-1932] P5: 75*[-957] P6: 4*[-3996] P7: 0*[]
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: OOB
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: bcnt=7val=5 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Sync: -3996 -> SyncFact: -0.25, Clock: NULL, Tol: 188, PattLen: 8 (2), Pulse: -940, -948, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050405050504040504040404040505050505050505060404050404050504040505050505050504050505040405040404040405050505050505050604040504040505040405050505050505040505050404050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[528] P2: 0*[-480] P3: 0*[-9096] P4: 48*[-1932] P5: 75*[-948] P6: 4*[-3996] P7: 0*[]
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: 77147 mb f a t proc 
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Sync: -3996 -> SyncFact: -0.25, Clock: NULL, Tol: 188, PattLen: 8 (2), Pulse: 516, -948, mStart: 0, MCD: 0, mtrunc: 1
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Signal: 06040405040405050404050505050505050405050504040504040404040505050505050505060404050404050504040505050505050504050505040405040404040405050505050505050604040504040505040405050505050505040505050404050404040404050505050505050506040405040405050404050505050505.  [254]
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 127*[-8] P1: 0*[528] P2: 0*[-480] P3: 0*[-9096] P4: 48*[-1932] P5: 75*[-948] P6: 4*[-3996] P7: 0*[]
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: OOB
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: bcnt=7val=1 mstart=0 mend=73 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.10 21:56:52 4: Arduino_Nano_433Mhz/msg READ: Sync: -3996 -> SyncFact: -0.25, Clock: NULL, Tol: 103, PattLen: 8 (2), Pulse: 516, -948, mStart: 0, MCD: 0, mtrunc: 1
[... cut ...]
2017.10.10 21:56:55 4: Arduino_Nano_868Mhz/msg READ: MS;P1=360;P2=-431;P4=552;P5=-637;P6=-9150;D=1612121212121212121212121245124512451245124512124512451212121212451245121245451212121212451212124512124545121212451245;CP=1;SP=6;R=0;O;
2017.10.10 21:56:55 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-435;P1=352;P2=556;P3=-640;D=010101023102310231023102310102310231010101010231023232310101010102310101023101023231010102310231;CP=1;R=250;
2017.10.10 21:56:55 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate

sidey79, 2017-10-10:
Du hast noch mal ein abweichendes Fehlerbild....
Da sind alle Stricke gerissen, im pattern Puffer steht nur noch schrott.
Der ist duch alle Abfragen durch und am Ende schlägt dann das hinzufügen fehl, da der message Puffer wirklich voll ist.
Ich könnte in so einem fall einfach ein Reset einbauen, was sicherlich nicht völlig verkehrt ist.

Zuerst habe ich mich aber entschieden, möchte ich feststellen, wo hier die Fehler mit dem pattern Puffer aufkommen. Das ist so gefühlt eine der Ursachen. Wie es halt so ist, zeigt sich der Fehler gerade im Moment bei mir nicht mehr :(

HomeAutoUser, 2017-10-10:
Ich habe sie wieder drin und lasse sie laufen.
Einen Zusammenhang wann es dazu kommt sehe ich bei mir noch nicht. Manchmal ist es sofort wenn ich den Empfänger flashte aber manchmal auch erst Minuten bzw. Stunden später.

Bei irgendwelchen "Zusammentreffen an Ereignissen" oder Signalen tritt es auf. Ursache finden wir!

HomeAutoUser, 2017-10-11:
Diesmal nach 7Stunden.

>2017.10.11 06:30:16 4: Arduino_Nano_868Mhz: Found manchester Protocol id 52 clock 490 RSSI -41.5 -> OS_PIR
2017.10.11 06:30:16 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1007;LH=950;SL=-524;SH=447;D=AE29974A2E59A97;C=487;L=60;R=50;
2017.10.11 06:30:16 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 487 RSSI -49 -> tfa 30.3208.0 
2017.10.11 06:30:16 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1009;LH=949;SL=-524;SH=447;D=ACD16B6218289D5;C=488;L=60;R=50;
2017.10.11 06:30:16 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 488 RSSI -49 -> tfa 30.3208.0 
2017.10.11 06:30:18 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1007;LH=950;SL=-524;SH=450;D=18289D5;C=488;L=28;R=204;
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-8975;P1=487;P2=-4048;P3=-2019;D=10121213121313131213121213121213131212131313131313121313131313131312121213;CP=1;SP=0;R=44;O;m2;
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 0 dmsg sD16CC080E000 length 40 RSSI = -52
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Matched MS Protocol id 68 -> PFR-130
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 68 dmsg sD16CC080E000 length 40 RSSI = -52
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz Dispatch: sD16CC080E000, Dropped due to short time or equal msg
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: 24781228 mb f a t proc 
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 95, PattLen: 8 (0), Pulse: -4032, 476, mStart: 0, MCD: 0, mtrunc: 1
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-8975;P1=483;P2=-4050;P3=-2022;D=10121213121313131213121213121213131212131313121313121213131212131313131313121313131313131312121213;CP=1;SP=0;R=44;m1;
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-8975;P1=483;P2=-4051;P3=-2022;D=10121213121313131213121213121213131212131313131313121313131313131312121213;CP=1;SP=0;R=44;
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Matched MS Protocol id 0 -> weather1
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 0 dmsg sD16CC080E000 length 40 RSSI = -52
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz Dispatch: sD16CC080E000, Dropped due to short time or equal msg
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Matched MS Protocol id 68 -> PFR-130
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz: Decoded MS Protocol id 68 dmsg sD16CC080E000 length 40 RSSI = -52
2017.10.11 06:30:22 4: Arduino_Nano_433Mhz Dispatch: sD16CC080E000, Dropped due to short time or equal msg
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Signal: 23252323262326262623262323262323262623232626262626262326262626262626232323262523232623262626232623232623232626232326262626262623262626262626262323232625232326262626262323232625232326232626262326232326232326262323262626262626232626262626262633736573736367.  [254]
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 0*[1202] P1: 0*[252] P2: 120*[-67] P3: 51*[-4050] P4: 0*[] P5: 5*[-9056] P6: 74*[-2024] P7: 4*[476]
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: bcnt=7val=3 mstart=0 mend=0 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 806, PattLen: 8 (0), Pulse: -4032, 476, mStart: 0, MCD: 0, mtrunc: 1
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Signal: 23252323262326262623262323262323262623232626262626262326262626262626232323262523232623262626232623232623232626232326262626262623262626262626262323232625232326262626262323232625232326232626262326232326232326262323262626262626232626262626262633736573736367.  [254]
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 0*[1202] P1: 0*[252] P2: 120*[-67] P3: 51*[-4041] P4: 0*[] P5: 5*[-9056] P6: 74*[-2024] P7: 4*[476]
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: 24781410 mb f a t proc 
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 806, PattLen: 8 (0), Pulse: 500, 476, mStart: 0, MCD: 0, mtrunc: 1
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Signal: 23252323262326262623262323262323262623232626262626262326262626262626232323262523232623262626232623232623232626232326262626262623262626262626262323232625232326262626262323232625232326232626262326232326232326262323262626262626232626262626262633736573736367.  [254]
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Pattern:  P0: 0*[1202] P1: 0*[252] P2: 120*[-67] P3: 51*[-4041] P4: 0*[] P5: 5*[-9056] P6: 74*[-2024] P7: 4*[476]
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: bcnt=7val=7 mstart=0 mend=0 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 state=0 addData overflow!!
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: 
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Sync:  Clock: NULL, Tol: 100, PattLen: 8 (0), Pulse: 500, 488, mStart: 0, MCD: 0, mtrunc: 1
2017.10.11 06:30:22 4: Arduino_Nano_868Mhz/msg READ: Signal: 23252323262326262623262323262323262623232626262626262326262626262626232323262523232623262626232623232623232626232326262626262623262626262626262323232625232326262626262323232625232326232626262326232326232326262323262626262626232626262626262633736573736367.  [254]
[... cut ...]
2017.10.11 06:47:23 4: Arduino_Nano_433Mhz/msg READ: Signal: 45555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555563030303030363036363636303036637600003330633036633666003060603000066633030363636003336030003600663630030.  [254]
2017.10.11 06:47:23 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 36*[-962] P1: 0*[913] P2: 0*[-540] P3: 39*[507] P4: 1*[92] P5: 149*[-32001] P6: 28*[-1948] P7: 1*[-3894]
2017.10.11 06:47:23 4: Arduino_Nano_433Mhz/msg READ: 25877432 mb f a t proc 
2017.10.11 06:47:23 4: Arduino_Nano_433Mhz/msg READ: 
2017.10.11 06:47:56 4: Arduino_Nano_433Mhz/keepalive ok, retry = 0

HomeAutoUser, 2017-10-12:
Ich fahre nun mal Tests, wo ich die einzelnen Signaltypen MU , MS , MC abstelle und dann laufen lasse.
Vielleicht ergeben sich darauf Erkenntnisse.

[49 more replies]

closed 2017-11-02 by sidey79