RFD-FHEM/SIGNALDuino #65: Firmware dev-r332_cc1101

Ralf9, 2017-10-06:
Hier soll es um Ideen, Vorschläge und todos für die Firmware dev-r332_cc1101 gehen.
- addData overflow
- Not detected as MS z.B. https://github.com/RFD-FHEM/SIGNALDuino/issues/38
- MU-Nachrichten werden z.T. als MC erkannt
- bei MU-Nachrichten Wiederholungen trennen

@sidey79 kannst Du bitte von der dev-r33_cc1101 einen neuen Branch dev-r332_cc1101 erstellen.
Ich denke es macht Sinn damit zu warten bis die Probleme in der dev-r33_cc1101 die die Abstürze verursachen behoben sind.

Ich sehe momentan nur die Trennung der Wiederholungen als wichtig an.
Der "addData overflow" und "MU-Nachrichten werden z.T. als MC erkannt" dürfte nicht so dringend sein.
Da gibt es z.Zt, noch einge todos mit höherer Priorität.

> Ein Ansatz bei MU Nachrichten zu trennen wäre, wenn wir einen low Pegel finden der länger als die anderen ist und vielleicht maximal 3x vorkommt. Dann könnte man diesen verwenden.
> Länger wäre dann vermutlich mindestens 2* maximal 10* ?

Evtl ist ein Ansatz, wenn wir eine Variable MuCut einführen, in der die mindestlänge einer Pause als Vielfaches der Clock gespeichert wird. z.B. bedeutet 10: Pause = 10 x clock.
Mit z.B. dem command GSMUCUT kann man es konfigurierbar machen und im EEPROM speichern.
Bei 10 wird dann nach einer Pause von mindestens 10 * clock gesucht

sidey79, 2017-10-06:
@Ralf9 

Abstürze habe ich gestern behoben... glaube ich :) => 86537af69be30ddc023419789201acf2a7cfa9e6
addData overflow hatte ich jetzt auch keine mehr.

Nach einigen Analysen habe ich noch festgestellt, dass ein BufferMove auf den kompletten Puffer nicht durchgeführt wird (macht auch keinen Sinn). => 98a2ed39667d7c496a596ca66d4732e9e717a1a3

Fehlererkennung MC Nachrichten habe ich bereits bearbeitet => 86537af69be30ddc023419789201acf2a7cfa9e6

Die Variabe MuCut würde ich vermutlich nicht wählen. Wieso soll das ein Anwender konfigurieren?
Das sollte doch eher möglichst immer funktionieren. Das ganze auf die den erkannten clockpulse auslegen finde ich gut.

Den teil aus getSync könnte man fast 1:1 beibehalten.
`
		for (int8_t p = patternLen - 1; p >= 0; --p)  // Schleife fuer langen Syncpuls
			uint16_t syncabs = abs(pattern[p]);
			if ((pattern[p] < 0) &&
				(syncabs < syncMaxMicros && syncabs / pattern[clock] <= syncMaxFact) &&
				(syncabs > syncMinFact*pattern[clock]) &&
				(histo[p] < messageLen*0.08) && (histo[p] > 1)
				)
			{
`

Was nicht aufgeht ist der teil in dem c berechnet wird: c wäre nach Ende der Schleife   identisch mit 
 messageLen-1
So wissen wir nicht, ob wir den richtigen Puls gerade bearbeiten.
`
				while (c < messageLen - 1)	
				{
					if (message[c + 1] == p && message[c] == clock) break;
					c++;
				}
`

Meine Ideen gehen wie folgt weiter:

1. Aufsteigendes Sortieren der Pulse (habe ich im MC Decoder schon mal so in der Art programmiert)
2. Beginnend mit dem längsten Low Puls, die obige Abfrage durchführen
3. Wenn der Abstand zwischen Beginn und dem 1. Vorkommen des low pegels > minbitlength ist, dann haben wir etwas brauchbares gefunden.

Wenn der Puls nur 1x vorkommt, naja dann geben wir halt alles aus, was vor diesem Puls ist und machen ein move. Den Puls selber würde ich im Puffer behalten.

HomeAutoUser, 2017-10-07:
@sidey79,

bitte nich zu früh freuen! Soeben habe ich deine neue FW getestet und schon erhalte ich erneut

> 2017.10.07 12:25:10 4: Arduino_Nano_868Mhz/msg READ: MU;P0=-23872;P1=411;P2=-385;P3=596;P4=-592;D=0121212121212121212121212343412121234343412123434341212341212123434343434121234121212121212123412341234343434121234341;CP=1;
2017.10.07 12:25:10 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: Starting demodulation at Position 1
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: applying postDemodulation
2017.10.07 12:25:10 4: Arduino_Nano_868Mhz: FHT80TF - door/window switch protolength 45
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: modified value after postDemodulation: 1 0 0 0 1 1 1 0 1 1 1 0 0 1 0 0 1 1 1 1 1 0 0 1 0 0 0 0 0 0 1 0
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: dispatching bits: 1 0 0 0 1 1 1 0 1 1 1 0 0 1 0 0 1 1 1 1 1 0 0 1 0 0 0 0 0 0 1 0
2017.10.07 12:25:10 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 70 dmsg T8EE4F902 length 32
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz Dispatch: T8EE4F902, test ungleich: disabled
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz Dispatch: T8EE4F902,  dispatch
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: dispatch T8EE4F902
2017.10.07 12:25:10 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: Starting demodulation at Position 1
2017.10.07 12:25:10 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: Starting demodulation at Position 1
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: applying postDemodulation
2017.10.07 12:25:10 5: Arduino_Nano_868Mhz: modified value after postDemodulation: 
2017.10.07 12:25:10 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-598;LH=599;SL=-397;SH=397;D=FFAF57577AABB7EEAAEC;C=331;L=78;
2017.10.07 12:25:22 4: Arduino_Nano_868Mhz/msg READ: val=2 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.10.07 12:25:22 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.10.07 12:25:22 4: Arduino_Nano_868Mhz/msg READ: val=2 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.10.07 12:25:22 4: Arduino_Nano_868Mhz/msg READ: val=3 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.10.07 12:25:22 4: Arduino_Nano_868Mhz/msg READ: val=4 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.10.07 12:25:22 4: Arduino_Nano_868Mhz/msg READ: val=1 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!
2017.10.07 12:25:22 4: Arduino_Nano_868Mhz/msg READ: val=2 msglen=254 bytecnt=126 valcnt=254 mTrunc=1 addData overflow!!

An der Erkennung der Protokolle wurd auch etwas verändert, so das die bisherige Erkennung der Entwicklerprotokolle FS20/FHT/FHT80b nicht mehr empfangen werden nach diesem Umstieg.
Vielleicht ist es die [Fehlerbehebung MC ](https://github.com/RFD-FHEM/SIGNALDuino/commit/86537af69be30ddc023419789201acf2a7cfa9e6).

HomeAutoUser, 2017-10-07:
@sidey79 zu meinem vorherigen Post,
1) DataOverflow ist noch bestehend.
2) RSSI Werte werden nicht mehr angezeigt. Meine Tabelle welche unverändert blieb schreibt die RSSI Werte des Empfängers nicht mehr mit. (+Arduino_Nano_868Mhz_RSSI - der Empfänger hat auch keine RSSI Ausgabe mehr beim empfangen Paket, also Empfänger Internals)
3) MC - MU Erkennung ist vorhanden bei Signalen welchen einen Overflow erzeugen.

sidey79, 2017-10-07:
Schade, wäre zu schön gewesen. :(

RSSI Werte werden nur noch übergeben, wenn ein CC1101 angebunden ist.

HomeAutoUser, 2017-10-07:
Ich VERWENDE einen CC1101!

> RSSI Werte werden nur noch übergeben, wenn ein CC1101 angebunden ist.

Ralf9, 2017-10-07:
evtl habe ich hier einen Denkfehler gemacht?

```
	#ifdef CMP_CC1101
			MSG_PRINT("R");  MSG_PRINT(rssiValue, HEX);  MSG_PRINT(SERIAL_DELIMITER);
	#endif	
```
muß dazu das 
#define CMP_CC1101 
zusätzlich noch in die signalDecoder.h geschrieben werden oder gibt es dafür eine eleganterere Lösung?

sidey79, 2017-10-07:
Die ganzen Konfig Parameter, müsste man in eine eigene Header Datei packen und dann in jeder bib einbinden.

Die Bibliothek wird ja unabhängig vom Hauptprogramm compilieren und dann nur gelinkt.

Gleiches Problem gibt es ja auch mit Output.h

sidey79, 2017-10-07:
@HomeAutoUser 

Ich habe verstanden, Du hast zwei SIGNALduinos... Wenn möglich, wäre es top wenn Du beide auf die gleiche Frequenz einstellst und dann einen mit MC Decoder aktiv und einen MC Devoder inaktiv laufen lässt. Mit Verbose 4 und aktiviertem mseclog würde mir das helfen.

HomeAutoUser, 2017-10-07:
@sidey79 ,
das machen wir sobald du das OK gegeben hast das die [Ausgabe mit addData](https://github.com/RFD-FHEM/SIGNALDuino/issues/62#issuecomment-334963733) reicht. So gehen wir das Schritt für Schritt an.

Ralf9, 2017-10-08:
> Die Variabe MuCut würde ich vermutlich nicht wählen. Wieso soll das ein Anwender konfigurieren?
> Das sollte doch eher möglichst immer funktionieren. Das ganze auf die den erkannten clockpulse auslegen finde ich gut.

Heisst das, daß Du davon ausgehst, daß dies so problemlos bei allen Protokollen funktionieren wird.
Mit der Variabe MuCut könnte man es bei Bedarf auch deaktiveren (=0 oder =255)

> Meine Ideen gehen wie folgt weiter:
>     Aufsteigendes Sortieren der Pulse (habe ich im MC Decoder schon mal so in der Art programmiert)
>     Beginnend mit dem längsten Low Puls, die obige Abfrage durchführen

und dann ggf runtergehen bis ca 9-10 mal clock und dann ein paar Werte davor abschneiden.
evtl reicht es erst ab minbitlength zu suchen

Zu den sync bei MS-Nachrichten,
müssten dies nicht auch MS-sync sein?
```
name		'AK-HD-4',	
id          	'59',
clockabs     	230, 						
zero		[-4,1],
one		[-1,4],
start		[-1,37],

name            'Clarus_Switch',
id              '62',
one             [3,-1],
zero            [1,-3],
start           [1,-35], # ca 30-40
clockabs        189, 

name            'Homeeasy',
id              '65',
one             [1,-5],
zero            [1,-1],
start           [1,-40],
clockabs        250,
```

Ralf9, 2017-10-08:
@HomeAutoUser 
> An der Erkennung der Protokolle wurd auch etwas verändert, so das die bisherige Erkennung der Entwicklerprotokolle FS20/FHT/FHT80b nicht mehr empfangen werden nach diesem Umstieg.
> Vielleicht ist es die Fehlerbehebung MC .

Funktioniert es noch wenn MC deaktiviert ist?
Dies wäre dann ja erstmal ausreichend, da momentan ein aktivertes MC bei 868MHz keinen Sinn macht.

Ralf9, 2017-10-28:
@HomeAutoUser
Ich versuche gerade das Trennen von Wiederholungen bei MU-Nachrichten einzubauen.
Für mich wären einige MU-Raw Nachrichten (mit allen Wiederholungen) interessant die Wiederholungen enthalten.

elektron-bbs, 2017-10-28:
Ich antworte mal stellvertretend. Ich habe mir heute noch einen Signalduino mit einem Arduino mini pro und einem CC1100 zusammen gesteckt. Firmware ist die aktuelle vom Do 26.10.2017. Wenn ich MC eingeschaltet habe, bekomme ich immer Ausgaben in der Art:

2017.10.28 20:50:13 4: sduino868/msg READ: MC;P0=-864;P1=393;P2=-395;P3=590;P4=-598;D=51212121212121212121212123412341234121234123412121234341212343412121212121212121234123412341234121212121212121212121210;O;
2017.10.28 20:50:13 4: sduino868/msg READ: MC;LL=-598;LH=590;SL=-395;SH=393;D=FFFB6EDEBAFFDB6FFFDF6;C=329;L=83;R=40;
2017.10.28 20:50:13 4: sduino868/msg READredu: MU;P1=396;P2=-394;P3=587;P4=-593;P5=-9304;D=151212121212121212121212123412341234121234123412121234341212343412121212121212121234123412341234121212121212121212121212123412121212341210;CP=1;R=40;

2017.10.28 20:52:08 4: sduino868/msg READ: MC;P0=-13472;P1=386;P2=-405;P3=602;P4=-588;D=51212121212121212121212123412341234121234123412121234341212343412121212121212121234123412341234121212121212121212121210;O;
2017.10.28 20:52:08 4: sduino868/msg READ: MC;LL=-588;LH=602;SL=-405;SH=386;D=FFFB6EDEBAFFDB6FFFDF6;C=330;L=83;R=41;
2017.10.28 20:52:08 4: sduino868/msg READredu: MU;P1=400;P2=-393;P3=602;P4=-591;P5=-9320;D=151212121212121212121212123412341234121234123412121234341212343412121212121212121234123412341234121212121212121212121212123412121212341210;CP=1;R=41;

Bei abgeschalteter MC-Erkennung sehen die Ausgaben folgendermaßen aus:

2017-10-28_20:21:20 FHT_5219 actuator: 0%
2017.10.28 20:21:20 4: sduino868/msg READredu: MU;P0=-29008;P1=397;P2=-387;P3=604;P4=-587;P5=-9304;D=0121212121212121212121212341234123412123412341212123434121234341212121212121212123412341234123412121212121212121212121212341212121234121512121212121212121212121234123412341212341234121212343412123434121212121212121212341234123412341212121212121212121212;CP=1;R=41;O;
2017.10.28 20:21:21 4: sduino868/msg READredu: MU;P0=-32001;P1=585;P2=-419;P3=395;P6=-622;D=012323232323232323232323216321632163232163216323232161632321616321632163232161632321616163216161632323232163232163232321632163232161632;CP=3;R=84;

2017-10-28_20:23:16 FHT_5219 actuator: 0%
2017.10.28 20:23:16 4: sduino868/msg READredu: MU;P0=-32001;P1=407;P2=-393;P3=592;P4=-593;P5=-9296;D=0121212121212121212121212341234123412123412341212123434121234341212121212121212123412341234123412121212121212121212121212341212121234121512121212121212121212121234123412341212341234121212343412123434121212121212121212341234123412341212121212121212121212;CP=1;R=39;O;
2017.10.28 20:23:16 4: sduino868/msg READredu: MU;P0=-32001;P1=578;P2=-416;P3=413;P6=-617;P7=312;D=012323232323232323232323216321632163232163216323232161632321616321632163232161632321616163216161632323232163232167232321632163232161632;CP=3;R=84;

2017-10-28_20:25:11 FHT_5219 actuator: 0%
2017.10.28 20:25:11 4: sduino868/msg READredu: MU;P0=-23104;P1=396;P2=-386;P4=592;P5=-598;P6=-9312;D=0121212121212121212121212451245124512124512451212124545121245451212121212121212124512451245124512121212121212121212121212451212121245121612121212121212121212121245124512451212451245121212454512124545121212121212121212451245124512451212121212121212121212;CP=1;R=39;O;
2017.10.28 20:25:12 4: sduino868/msg READredu: MU;P0=-32001;P1=586;P2=-411;P3=396;P6=-610;D=012323232323232323232323216321632163232163216323232161632321616321632163232161632321616163216161632323232163232163232321632163232161632;CP=3;R=84;

2017-10-28_20:27:07 FHT_5219 actuator: 0%
2017.10.28 20:27:07 4: sduino868/msg READredu: MU;P0=-4664;P1=405;P2=-396;P3=600;P4=-585;P5=-9296;D=0121212121212121212121212341234123412123412341212123434121234341212121212121212123412341234123412121212121212121212121212341212121234121512121212121212121212121234123412341212341234121212343412123434121212121212121212341234123412341212121212121212121212;CP=1;R=39;O;
2017.10.28 20:27:09 4: sduino868/msg READredu: MU;P0=-32001;P1=579;P2=-397;P3=401;P6=-621;D=012123232121232123232323216321632163232163216323232161632321616321632163232161632321616163216161632323232163232163232321632163232161632;CP=3;R=83;

2017-10-28_20:29:02 FHT_5219 actuator: 0%
2017.10.28 20:29:02 4: sduino868/msg READredu: MU;P0=-17240;P1=399;P2=-380;P3=586;P4=-601;P5=-9296;D=0121212121212121212121212341234123412123412341212123434121234341212121212121212123412341234123412121212121212121212121212341212121234121512121212121212121212121234123412341212341234121212343412123434121212121212121212341234123412341212121212121212121212;CP=1;R=41;O;
2017.10.28 20:29:03 4: sduino868/msg READredu: MU;P0=-32001;P1=608;P2=-401;P3=412;P5=-584;D=012323232323232323232323215321532153232153215323232151532321515321532153232151532321515153215151532323232153232153232321532153232151532;CP=3;R=84;

2017-10-28_20:38:40 FHT_5219 actuator: 0%
2017.10.28 20:38:40 4: sduino868/msg READredu: MU;P0=-32001;P1=396;P2=-385;P3=593;P4=-589;P5=-9296;D=0121212121212121212121212341234123412123412341212123434121234341212121212121212123412341234123412121212121212121212121212341212121234121512121212121212121212121234123412341212341234121212343412123434121212121212121212341234123412341212121212121212121212;CP=1;R=41;O;
2017.10.28 20:38:41 4: sduino868/msg READredu: MU;P0=-32001;P1=403;P2=-411;P5=591;P6=-618;D=012121212121212121212121256125612561212561256121212565612125656125612561212565612125656561256565612121212561212561212125612561212565612;CP=1;R=84;

[... cut ...]
http://fhz4linux.info/tiki-index.php?page=FHT%20protocol bzw. http://fhz4linux.info/tiki-index.php?page=FS20%20Protocol oder auch http://www.cc2net.de/Foren/CC2Net_Forum/lesen.php?eintrag=8457

Ansonsten funktioniert diese Firmware-Version auf einem Radino-CC1101 auch mit eingeschalteten MC bei mir tadellos. Ich hatte nur Probleme alle Debug-Ausgaben auszuschalten, damit das Programm in den Speicher passt. Ich empfange damit aktuell folgende Protokolle: WS2000, CUL_TCM97001 W174, CUL_TX WS7035 WS7053, SD_WS07 und SD_WS37.

sidey79, 2017-10-30:
Hi @elektron-bbs 

Ich habe mir mal
```
MC;P0=-864;P1=393;P2=-395;P3=590;P4=-598;D=51212121212121212121212123412341234121234123412121234341212343412121212121212121234123412341234121212121212121212121210;O;�
2017.10.28 20:50:13 4: sduino868/msg READ: �MC;LL=-598;LH=590;SL=-395;SH=393;D=FFFB6EDEBAFFDB6FFFDF6;C=329;L=83;R=40;�
```

Angesehen. Aktuell weiss ich nicht, wieso man das nicht als MC erkennen sollte.
Es passt einfach alles.

Ralf9, 2017-10-30:
Das Verhältnis von S (400) zu L (600) ist 1.5, dies müsste doch bei MC 2 sein?

sidey79, 2017-10-30:
Naja, genau 2 ist es meist nie.. 1,5 bis 2,5 ist so im Bereich...

Ralf9, 2017-10-30:
Kann es bei MC Nachrichten einen sehr langen Puls (hier P5=-9304) ungefähr in der Mitte der Nachricht geben?

MU;P0=-29008;P1=397;P2=-387;P3=604;P4=-587;P5=-9304;D=0121212121212121212121212341234123412123412341212123434121234341212121212121212123412341234123412121212121212121212121212341212121234121**5**12121212121212121212121234123412341212341234121212343412123434121212121212121212341234123412341212121212121212121212;CP=1;R=41;O;

sidey79, 2017-10-30:
Klar, das kann schon sein.

elektron-bbs, 2017-10-31:
Hallo @sidey79 

> Angesehen. Aktuell weiss ich nicht, wieso man das nicht als MC erkennen sollte.
 Es passt einfach alles.

Das sehe ich anders. Es passt schon mal nicht, das der Signalduino eine Nachricht in 3 verschiedenen Varianten ausgibt:

2017.10.28 20:50:13 4: sduino868/msg READ: �MC;P0=-864;P1=393;P2=-395;P3=590;P4=-598;D=51212121212121212121212123412341234121234123412121234341212343412121212121212121234123412341234121212121212121212121210;O;�
2017.10.28 20:50:13 4: sduino868/msg READ: �MC;LL=-598;LH=590;SL=-395;SH=393;D=FFFB6EDEBAFFDB6FFFDF6;C=329;L=83;R=40;�
2017.10.28 20:50:13 4: sduino868/msg READredu: MU;P1=396;P2=-394;P3=587;P4=-593;P5=-9304;D=151212121212121212121212123412341234121234123412121234341212343412121212121212121234123412341234121212121212121212121212123412121212341210;CP=1;R=40;

Die übergebene Anzahl Bits ist auch unschlüssig.
Die erste MC Ausgabe ist wohl die unvollständige Wiederholung nach der Pause von 9 mSek (erkennbar an der 5 als erstes).
Die 2. MC ist offensichtlich der erste Teil komplett und ein Teil der Wiederholung (Beginn jeweils mit FFF als Präambel, sollte aber eigentlich 000 sein).
Die 3. Ausgabe als MU passt von der Länge (67 Bit) und ist aber die Wiederholung der Nachricht (erkennbar am Beginn mit 9 mSek Pause).

sidey79, 2017-10-31:
Ich habe eine Debug MC Ausgabe aktiviert, damit ich Mal genauer sehe, was im Nachrichten Puffer ist, wenn es als MC erkannt wird.

Davon nicht beirren lassen. Das was da drin steht, ist valides Manchester. Wenn Du da anderer Meinung bist, dann gibt mit doch bitte genauere Hinweise.

sidey79, 2017-10-31:
Habe mir das mit MU ausgegebene noch mal genauer angesehen, wieso das nicht als MC erkannt wird.
Es ist eine Toleranzfrage, ließe sich so hinbiegen, dass es als MC erkannt wird.

Ist halt die Grundsatzfrage, ist es MC, weil es zufällig passt oder eben nicht?

elektron-bbs, 2017-10-31:
Ich denke nicht, das das Manchester ist. Zitat: "Manchester Code wird durch Anwenden der XOR Operation auf das Taktsignal und den Datenstrom (NRZ-Signal) erzeugt." Das bedeutet für mich, das ein Flankenwechsel immer im gleichen Takt erfolgen muss und nicht mal nach 400 µS und mal nach 600 µS. Vielleicht hast du ja die Toleranz für Manchester-Erkennung zu großzügig eingestellt.

sidey79, 2017-10-31:
Ja, die Toleranz für einen Takt ist 0,5 und für zwei Taktlängen 0,8 . Das ist leider notwendig, damit die Messungenauigkeiten sowohl beim Senden als auch beim Empfangen kompensiert werden können. Ich könnte die Toleranz für eine Taktlänge auf 0,4 reduzieren. Irgendwo wird das halt vermutlich wieder ein neues Problem aufwerfen.

elektron-bbs, 2017-11-01:
Wenn ich jetzt die richtige Stelle gefunden habe:

			if (pdec->inTol(clockpulse, abs(aktpulse), clockpulse*0.5))
				pshort = true;
			else if (pdec->inTol(clockpulse*2, abs(aktpulse), clockpulse*0.80))
				plong = true;

würde ich doch eher dazu tendieren, für plong die Toleranz zu verringern. Wenn ich richtig rechne, passt bei dieser Toleranz bei einem clockpulse von 400 ein aktpulse von 600 natürlich auch. Erst wenn ich einen Multiplikator von 0.4 verwende, wird 600 nicht mehr gültig sein. Ich habe jetzt probehalber beide Multiplikatoren auf 0.4 gesetzt. Dann werden die FHT und FS20 nicht mehr als MC erkannt. Mangels Sensoren mit MC-Kodierung kann ich allerdings keine Gegenprobe machen.

Ralf9, 2017-11-01:
Wenn Du die Toleranz änderst, müsstest Du prüfen ob dann die anderen MC Protocol Ids (10 11 12 12.1 18 43 47 52 57 58) noch funktionieren.
Mir ist noch nicht klar was dagegen spricht bei 868 MHz den MC-Decoder zu deaktivieren. Sind bei 868 MHz Sensoren bekannt die Manchester verwenden?

Ich würde es wichtiger finden, daß der MC-Decoder wieder so gut funktioniert wie vor diesen Anpassungen:
https://github.com/RFD-FHEM/SIGNALDuino/commits/dev-r33_fixmc
Bei den Hideki Sensoren sind die Daten nach der Dekodierung z.T. invertiert und z.T. nicht invertiert.
Der Maverick Grillthermometer funktioniert anscheinend inzwischen auch nicht mehr:
https://forum.fhem.de/index.php/topic,77756.msg697123.html#msg697123
> Mit dem Signalduino empfange ich auch nur noch Unsinn.
> (Das hat einmal auch funktioniert!)

sidey79, 2017-11-01:
Mir ist ehrlich gesagt nicht klar, wieso wir die Probleme aus #62 nun hier mit vermischen.
Was wie zusammen hängt wird immer unklarer. Der Zusätzliche Branch macht es auch nicht leichter, da er 12 commits zurück hängt.

Nun noch mal was generelles zum Thema Manchester:
Es gibt von Atmel eine recht gute Beschreibung inkl. Referenz Implementierung:
http://www.atmel.com/images/atmel-9164-manchester-coding-basics_application-note.pdf

Charakteristisch für Manchester:

- Gleichanteilsfreie Übertragung
- Takt lässt sich aus dem empfangenen Signal bestimmen

An dieser habe ich mich orientiert. Aufgrund diverser Hersteller hat der MC Decoder aber leider viele "Ergänzungen" erfahren. Die machen das ganze leider auch etwas komplexer.

In der Beschreibung wird eine Toleranz von bis zu  +-50% empfohlen und im Beispiel auch umgesetzt:
Für short:
`if((tmp >= (clkT*0.5)) && (tmp <= (clkT*1.5)))`

Für long:
`if((tmp >= (clkT*1.5)) && (tmp <= (clkT*2.5)))`

Der MC Decoder schnappt sich nicht, was mit mehreren Nullen beginnt. Er schnappt sich alles, wo die Prüfroutinen ein valides Manchester Signal erkennen. Bei dieser Übertragung passt das leider (manchmal).

Anpassungen der Toleranzen:
- Toleranz für die short Pegel auf 40% reduzieren. Für die long Pegel lassen wir es bei 80% des Taktes.

Ich könnte mir vorstellen, dass wir das Signal noch Zusätzlich auf die Signalfolgen prüfen und gegen die Anzahl der unterschiedlichen Sequenzen prüfen.  Also immer zwei Pulse ergeben dann eine Sequenz.
Haben wir genau zwei unterschiedliche Sequenzen, ist das für Manchester eher untypisch (abgesehen davon, wenn nur FFFFF übertragen werden würde).  EIn bisschen was habe ich dafür schon entwickelt, es läuft leider aber noch nicht rund.

sidey79, 2017-11-01:
Keines Update, bei einer Toleranz von nur 0,40 gibt es Probleme mit der Oregon V2 Dekodierung :(

HomeAutoUser, 2017-11-02:
Guten Morgen,
ich verfolge das Ganze ja auch im Hintergrund und mache mir diesbezüglich Gedanken.
Wir müssen natürlich aufpassen, das wir wirklich verschiedene Dinge nicht vermischen erstmal.
Da muss ich @sidey79 zustimmen.

> Mir ist ehrlich gesagt nicht klar, wieso wir die Probleme aus #62 nun hier mit vermischen.

Diesbezüglich würde ich einschätzen, das wir vermutlich an einem Punkt sind, wo wir dem Problem von [#62](https://github.com/RFD-FHEM/SIGNALDuino/issues/62) auf die Schlichte kamen, da es erstmal nicht mehr auftrat aber dafür der Empfang von manchen Sensoren nicht mehr gegeben war. Das ist verschuldet vermutlich Aufgrund des ineinander greifen.

>Nun noch mal was generelles zum Thema Manchester:

Das Thema ist auch sehr ausschlaggebend für das weitere Projekt finde ich. Natürlich ist es schwierig da eine klare Definition zu finden um alle 3 Protokolle MU - MC - MS zu unterscheiden. Dennoch sollte es Ziel sein, das man ohne Abschaltung von etwas zum Erfolg kommt.

@Ralf9 - Meinung
>Mir ist noch nicht klar was dagegen spricht bei 868 MHz den MC-Decoder zu deaktivieren. Sind bei 868 MHz Sensoren bekannt die Manchester verwenden?

Ich würde das vorgehen vom Abschalten nicht als Ursache wählen wollen. Das Thema der Erkennung wie o. angesprochen wird somit nur "vertagt" und zu einem späteren Zeitpunkt kommen wir wieder dort an den Punkt. Dennoch wissen wir nicht was auf dem 868 MHz Band noch an Sensoren kommen wird, denn dann wäre es verkehrt sich diese Option zu verbauen.

>Ich würde es wichtiger finden, daß der MC-Decoder wieder so gut funktioniert wie vor diesen Anpassungen:

Ja dem stimme ich auch zu, ABER under dem o.g. Aspekt, das man nicht einfach etwas abschaltet. Das Projekt kann auf verschiedenen Frequenzen benutzt werden und so sollten wir auch weiter denken. Es kann immer stets ertwas neues kommen was wir entdecken aber nicht immer wieder an den Punkt kommen sollten, das das Thema vertagt wird und bei neuen Sensoren aufgekaut wird. Zumal wenn man eine zuverlässige Definition findet OHNE das man willkürlich MC abschalten müsste, so verhindert wir auch USER Anfragen die vielleicht das selbige Problem haben. **Lasst uns bitte so denken, Fehler finden, korrigieren aber nicht vertagen.**

>Das Problem mit den fälschlich als MC deklarierten Protokollen hat ja nichts mit dem benutzten Frequenzband zu tun. Wir hatten das auch schon beim WS7000/WS2000 (RFD-FHEM/RFFHEM#136). Dort hat sich der Fehler mit dem Einbau der Änderungen aus dem dev-r33_fixmc erledigt. Nun taucht er wieder auf. Es wäre doch besser, eine richtige Lösung zu finden, als nur erst mal etwas abzuschalten.

Ich bin da ganz @elektron-bbs seiner Meinung.

>Bei den Hideki Sensoren sind die Daten nach der Dekodierung z.T. invertiert und z.T. nicht invertiert.
Der Maverick Grillthermometer funktioniert anscheinend inzwischen auch nicht mehr:

Das sind Dinge welchen wir nachgehen müssen aber dazu muss ein "klarer Grund" existieren" Wir können ungern etwas Universelles zum Teil spezifizieren.

Um nun auf die Punktliste aus dem ersten Faden zu kommen, @sidey79 wie weit kannst du einschätzen, das das Problem 
>addData overflow

behoben ist? Meine Einschätzung habe ich ja schon geschrieben. Auftreten war bisher nicht vorhanden (nicht mehr erhaltene Sensoren, erstmal ausgeklammert) Wenn du sagst, "fertig", dann bitte den code als Basis nehmen um den nächsten Punkt widmen. Da würde ich die "allgemeine Erkennung" der Sensoren erstmal anstreben. Danach dann die MC Erkennung auf allen Bändern (486/868) ...

Wir derzeit aktiven 4 Mitglieder @Ralf9 @sidey79 @elektron-bbs sind zusammen an einer Vielzahl an Sensoren wo man doch erstmal einen "klaren" Schnitt hinbekommen sollte.

elektron-bbs, 2017-11-02:
@sidey79 

> •Toleranz für die short Pegel auf 40% reduzieren. Für die long Pegel lassen wir es bei 80% des Taktes.

Ich habe jetzt mal folgende Varianten durchprobiert, dein Vorschlag bringt auch keine Verbesserung.

//pshort *0.4, plong *0.8 MU Erkennung geht nicht
//pshort *0.5, plong *0.5 MU Erkennung geht besser
//pshort *0.4, plong *0.4 MU Erkennung geht

Ist eigentlich auch logisch:
Bei gegebenen Pulszeiten von 400 und 600 bei FS20/FHT u.s.w., ergibt 400*0.5=600 - das passt dann manchmal, je nach Abweichung vom Soll.
Die Oregon V2 haben aber auch verdammt ähnliche Intervalle: 396, 884, 581, 1069 (laut Doku).

sidey79, 2017-11-02:
Tia, das ist durchaus etwas herausfordernd finde ich.

Die Prüfung isManchester kennt da nur schwarz oder weiß. Wenn ein Kriterium nicht passt, dann wird abgebrochen. Wie gut die passen oder nicht, das wird derzeit nicht berücksichtigt.
Vielleicht wäre das ein Ansatz etwas zu verändern.

Wenn ich mir das Signal optisch ansehe, dann fällt mir ja auf, dass es genau zwei Signalfolgen gibt und das ist für Manchester eher untypisch.

sidey79, 2017-11-02:
@HomeAutoUser @elektron-bbs 

Ich habe jetzt das mit den Sequenzen in die Firmware eingebaut. Bitte mal testen. Wenn das hilft gehen wir das mit den manchmal invertierten Signalen an :)

HomeAutoUser, 2017-11-03:
Guten Morgen,
@sidey79 ich habe nun deine FW mal auf den Empfänger geladen. Auf jedenfall muss ich immer noch MC deaktivieren. Hier mal dieAusgabe der falschen Erkennung.

>2017.11.03 07:06:36 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ:  seq+:101 seq+:104 seq+:154 seq+:154 
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ: MC;P0=-612;P4=378;P5=-409;P6=612;D=47454545454545454545454545604545456045606045604545456045606060454545454545454545456045604545606045454545454545454545456060456060606060604;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-612;LH=612;SL=-409;SH=378;D=7FD6AAA;C=335;L=27;R=212;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ:  seq+:147 seq+:145 seq+:145 seq+:145 
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ: MC;P0=-612;P4=378;P5=-409;P6=612;D=4;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-612;LH=612;SL=-409;SH=378;D=FFFBDADED5FFDBAFFF5AAA8;C=335;L=89;R=212;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ:  seq+:101 seq+:121 seq+:104 seq+:151 
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ: MC;P1=-613;P2=405;P6=-398;P7=611;D=2;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-613;LH=611;SL=-398;SH=405;D=6EBAAF77DAAA;C=337;L=47;R=219;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READ:  seq+:101 seq+:123 seq+:124 seq+:154Mu;���;���;���;���;d#$TTS$TTS!##$S#$S###$TTTTTQ#;C3;R48;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-412;P1=400;P2=260;P3=580;P4=-640;D=0101020341010103434343410103434343410103410341010341034343434341010;CP=1;R=72;
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.03 07:06:38 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.03 07:06:39 4: Arduino_Nano_868Mhz/msg READ:  seq+:123 seq+:143 seq+:153 seq+:153Mu;���;�Ղ;��;���;D#CSS@@P@@@P@@@P@@;C4;R48;
2017.11.03 07:06:39 4: Arduino_Nano_868Mhz/msg READ:  seq+:123 seq+:143 seq+:113 seq+:143Mu;���;��;��;���;D#CCCC``````@`@```@``;C6;R48;
2017.11.03 07:06:39 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-3384;P1=200;P2=-423;P3=587;P4=-594;P5=405;D=0123434343452523434525252525252523434343434345234343434;CP=5;R=72;
2017.11.03 07:06:39 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.03 07:06:39 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.03 07:06:39 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=396;P1=-451;P2=546;P3=-668;P5=-30808;P6=176;D=123230101232323230101012301012301012323232323230123230525610103230103010123232323010123230;CP=0;R=216;
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz/msg READ:  seq+:101 seq+:123 seq+:143 seq+:143Mu;���;���;�ۂ;���;d#CCE#CC%%'%#G%#E%%'%%#CE#CE;C5;R48;
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-31252;P1=637;P2=403;P3=-596;P4=-410;D=01023131313132424241313132413131324242424132424132424241324242424242413;CP=2;R=72;
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.03 07:06:40 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.03 07:06:41 4: Arduino_Nano_868Mhz/msg READ:  seq+:101 seq+:121 seq+:123 seq+:145Mu;�ς;�;���;���;d!#EEA!!%A!!%EEEA%EA%EEa%C#EEEA%;C5;R48;
2017.11.03 07:06:42 4: Arduino_Nano_868Mhz/msg READ:  seq+:101 seq+:121 seq+:103 seq+:123 
2017.11.03 07:06:42 4: Arduino_Nano_868Mhz/msg READ: MC;P0=-404;P1=399;P2=-601;P3=609;D=15101010101010101010101010321010321010101010321010321032321010321010101010101010103210321010323210101010101010101010103232323232323210321;
2017.11.03 07:06:42 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-601;LH=609;SL=-404;SH=399;D=1BAFFF55568;C=335;L=41;R=49;
2017.11.03 07:06:42 4: Arduino_Nano_868Mhz/msg READ:  seq+:115 seq+:110 seq+:132 seq+:132 
2017.11.03 07:06:42 4: Arduino_Nano_868Mhz/msg READ: MC;P0=-404;P1=399;P2=-601;P3=609;D=16;
2017.11.03 07:06:42 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-601;LH=609;SL=-404;SH=399;D=FFFBBF76BBFF6EBFFD555A;C=335;L=87;R=49;
2017.11.03 07:07:13 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=112;P1=-386;P2=621;P3=-598;P4=-8956;P5=410;D=0151515151515151235123515123235151515151515151515151515151512351512351545151515151515151515151512351515123512323235151235151515151512351515151515151515123512351512323515151515151515151515151515151235151235150;CP=5;R=9;
2017.11.03 07:07:13 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
[... cut ...]
2017.11.03 07:07:35 4: Arduino_Nano_868Mhz/msg READ:  seq+:101 seq+:123 seq+:103 seq+:125 
2017.11.03 07:07:35 4: Arduino_Nano_868Mhz/msg READ: MC;P0=431;P1=-398;P2=235;P3=-182;D=523212523032123032323;
2017.11.03 07:07:35 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-398;LH=431;SL=-182;SH=235;D=8A0578;C=207;L=21;R=199;
2017.11.03 07:07:36 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0

Desweiteren ist mir ebenso aufgefallen, die Hideki Erkennung ist schlechter.
Das sollten wir auch parallel du dem MC - Bug bewältigen. Hast du auch die Erkenntnis @Ralf9 mit der Hideki - Erkennung?

elektron-bbs, 2017-11-03:
Bei mir sieht es ähnlich aus :-(
[FHT_2017-11-03.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1441449/FHT_2017-11-03.log)
[FS20_2017-11-03.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1441451/FS20_2017-11-03.log)

sidey79, 2017-11-03:
Habe noch nicht alle daten verifiziert, aber es gibt schon mal ein kleines Update..

sidey79, 2017-11-03:
@elektron-bbs 
Also FHT, dürfte jetzt als MU erkannt werden. Habe es mit deinen Logs verifiziert

HomeAutoUser, 2017-11-04:
Guten Morgen,
@sidey79  schomal "daumenhoch" ;-)
Die FHT´s werden nun zuverlässiger erkennt. Ich habe dies soeben mal laufen lassen.
Für die FHTTK Kontakte gibt es noch verbesserung was die MU Erkennung angeht.

>2017.11.04 10:41:11 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-2104;P1=393;P2=-399;P3=583;P4=-608;D=012121212121212121212121234121212343412123434123412343412343434121234341212121212121212121212341234341234341212341212;CP=1;R=4;
2017.11.04 10:41:11 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.04 10:41:11 4: Arduino_Nano_868Mhz: FHT80TF - door/window switch protolength 45
2017.11.04 10:41:11 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 70 dmsg T195B3002 length 32 RSSI = -72
2017.11.04 10:41:11 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.04 10:41:12 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.04 10:41:12 1: PERL WARNING: Use of uninitialized value $returnvalues[0] in join or string at ./FHEM/00_SIGNALduino.pm line 4064.
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz/msg READ:  +seq[:41]
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=390;P2=-410;P3=588;P4=-595;D=012121212121212121212121234121212343412123434123412343412343434121234341212121212341212121212341212121234341212341234;CP=1;R=4;
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz: FHT80TF - door/window switch protolength 45
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 70 dmsg T195B3082 length 32 RSSI = -72
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.04 10:41:14 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READ:  +seq[:30]
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READ:  +seq[:103]
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READ:  +seq[:14]
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-619;P1=387;P2=-411;P3=598;D=0121230121212121212301212121212121212123012303012303012301212303012301212301230121230301230121412121212121212121212121230121212301230303012123012121212121230121212121212121212301230301230301230121230301230121230123012123030123012;CP=1;R=14;
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.04 10:41:18 1: PERL WARNING: Use of uninitialized value in addition (+) at ./FHEM/00_SIGNALduino.pm line 4236.
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz: FHT80 - roomthermostat protolength 54
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 73 dmsg 810c04xx0909a00117400000B634 length 48 RSSI = -67
2017.11.04 10:41:18 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:41]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:14]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:43]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:56]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ: MC;P1=394;P4=-398;P5=595;P6=-619;D=1714141414141414141414141456141414561456561456141414561456565614141414141414141414561456561456561456141414141414141414565656145656565;O;
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-619;LH=595;SL=-398;SH=394;D=BDADED5FFDAD6FFD5AAA;C=334;L=79;R=230;
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:17]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:71]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:14]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:41]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:56]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READ:  +seq[:61]
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=397;P4=-398;P5=593;P6=-619;D=171414141414141414141414145614141456145656145614141456145656561414141414141414141456145656145656145614141414141414141456565614565656565610;CP=1;R=230;
2017.11.04 10:41:43 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
[... cut ...]
2017.11.04 10:51:35 4: Arduino_Nano_868Mhz/msg READ:  +seq[:105]
2017.11.04 10:51:35 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.04 10:51:35 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.04 10:51:35 4: Arduino_Nano_868Mhz/msg READ:  +seq[:25]
2017.11.04 10:51:35 4: Arduino_Nano_868Mhz/msg READ:  +seq[:57]
2017.11.04 10:51:35 4: Arduino_Nano_868Mhz/msg READ: MC;P0=-398;P1=606;P2=-609;P5=395;D=57;
2017.11.04 10:51:35 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-609;LH=606;SL=-398;SH=395;D=FFFBDADED5FFDAD6FFD5AAA;C=334;L=91;R=234;

elektron-bbs, 2017-11-04:
Ja, mit dem FHT sieht das jetzt schon sehr gut aus. In ca. 1 h wurde nur eine Nachricht als MC definiert. Anbei die Logs vom SIGNALduino und CUL zum Vergleich. Ich habe den Regler zwischendurch verstellt, um verschiedene Werte in die Ausgabe zu bekommen. Bei dem Test wurde bei mir ein FHTTK fälschlicherweise angelegt. Da passte wohl zufällig die Prüfsumme. Die Sub für den FHTTK habe ich deshalb noch um eine Längenprüfung vervollständigt ( @HomeAutoUser müsste [das](https://github.com/RFD-FHEM/RFFHEM/issues/171) dann mal testen).
[FHT_2017-11-04.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1443534/FHT_2017-11-04.log)
[FHT_5219-2017-11-04_CUL.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1443535/FHT_5219-2017-11-04_CUL.log)

Bei FS20 funktioniert das noch nicht. Anbei auch nochmal einige Logausgaben:
[FS20_2017-11-04.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1443543/FS20_2017-11-04.log)

Was mir beim anschließenden Test mit abgeschaltetem MC auffiel: Die Nachrichten von FS20 werden neuerdings gesplittet. Hast du diesbezüglich auch schon etwas eingebaut, oder hat sich das zufällig ergeben?
[FS20_2017-11-04_split.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/1443545/FS20_2017-11-04_split.txt)

closed 2020-04-12 by sidey79