RFD-FHEM/SIGNALDuino #68: Überlegungen Fehlersuche / Debugging

HomeAutoUser, 2017-11-04:
Hallo, da wir ja gelegendlich auf der Suche nach Fehlern sind wieso Protokolle teilweise nach einem Bugfix oder einer Verbesserung schlechter erkannt werden, so machte ich mir mal Gedanken.

Gibt es eine Möglichkeit welche dazu beitragen kann sich der Sache sinnvoll anzunehmen?

Ich hatte die Überlegung 2 Empfänger welche einmal die gute und die "schlechtere" FW haben zu vergleichen. Zusätzlich dient eine Tabelle in FHEM welche mir das letzte Protokoll / DSMG anzeigt und den Empfangszeitpunkt mit Count. So sieht man, wenn Bsp ein Signal mit Wiederholungen von dem einen nur zum Teil erkannt wird. Aus dieser Erkenntnis würde ich im Logfile schauen wie die Daten aussehen welche verarbeitet werden. Wäre das Sinnvoll oder kann neune Erkenntnisse bringen?

Einziges "Problem " was ich sehe, das ggf. in den FW Versionen verschiedene Debugausgaben sind welche man nicht direkt vergleichen kann.

Gibt es diesbezüglich Ideen der herangehensweise?

Im Falle aktuellen Hideki-Protokoll ist es so, das die Nachrichten zu 2/3 schlechter erkannt werden wenn ich die FW @sidey79 SIGNALDuino-dev-r33_cc1101_171103 mit @Ralf9 SIGNALDuino-dev-r33_cc1101_171029_Ralf vergleiche.

HomeAutoUser, 2017-11-04:
Schaut Euch mal bitte diesen Abschnitt des Logfiles an.
Wenn ich richtig das richtig erkenne, werden bei dem Nano teilweise Nachrichten als MC definiert aber so wie dahinter die Definition geschrieben ist, kenne ich das "nur von" Mu bisher. Ist das ein "Bug" ?

>2017.11.04 11:35:03 4: radino_Stick_433Mhz/msg READ: MC;LL=-1013;LH=935;SL=-524;SH=449;D=1D668B5978C1732ABC1E08;C=486;L=85;R=29;
2017.11.04 11:35:03 4: radino_Stick_433Mhz: Found manchester Protocol id 12 clock 486 RSSI -59.5 -> Hideki protocol
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:101]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:12]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:21]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:13]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:34]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:43]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:42]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:24]

>2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ: MC;P1=946;P2=-1007;P3=-531;P4=456;D=0121213434243431212431342431213434212431212431342124343431343424313434343421243431342431342121212124343431343434342434343134343434243;

>2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1007;LH=946;SL=-531;SH=456;D=A8EB345ACBC60B9955E0F06;C=489;L=91;R=3;
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12 clock 489 RSSI -72.5 -> Hideki protocol
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz: hideki protocol converted to hex: 75CABACAE1BECC557C7C01 with 91 bits, messagestart 1
2017.11.04 11:35:03 4: Hideki_Parse Arduino_Nano_433Mhz incomming P12#75CABACAE1BECC557C7C01
2017.11.04 11:35:03 4: Hideki_Parse SensorTyp = 30 decodedString = 755ece5e23c254ff848403
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz decoded Hideki protocol model=Hideki_30, sensor id=5e, channel=2, bat=ok, temp=22.3, humidity=54
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:101]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:12]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:23]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:32]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:34]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:43]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:21]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:14]

>2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ: MC;P1=-1014;P2=452;P3=-526;P4=939;D=0123232343232123432323232141232343212343214123234123;

>2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1014;LH=939;SL=-526;SH=452;D=78C1732EC;C=488;L=34;R=46;
2017.11.04 11:35:03 4: radino_Stick_433Mhz/msg READ: MC;LL=-1007;LH=943;SL=-513;SH=460;D=A8EB345AEBC60B9955E1E74;C=487;L=90;R=30;
2017.11.04 11:35:03 4: radino_Stick_433Mhz: Found manchester Protocol id 12 clock 487 RSSI -59 -> Hideki protocol
2017.11.04 11:35:03 4: radino_Stick_433Mhz: hideki protocol converted to hex: 75CABA8AE1BECC553C4601 with 91 bits, messagestart 1
2017.11.04 11:35:03 4: Hideki_Parse radino_Stick_433Mhz incomming P12#75CABA8AE1BECC553C4601
2017.11.04 11:35:03 4: Hideki_Parse SensorTyp = 30 decodedString = 755ece9e23c254ff44ca03
2017.11.04 11:35:03 4: radino_Stick_433Mhz decoded Hideki protocol model=Hideki_30, sensor id=5e, channel=2, bat=ok, temp=22.3, humidity=54
2017.11.04 11:35:03 4: radino_Stick_433Mhz/msg READ: MC;LL=-1003;LH=953;SL=-522;SH=455;D=51D668B5B78C1732ABC0FA8;C=488;L=89;R=30;
2017.11.04 11:35:03 4: radino_Stick_433Mhz: Found manchester Protocol id 12 clock 488 RSSI -59 -> Hideki protocol
2017.11.04 11:35:03 4: radino_Stick_433Mhz: hideki protocol converted to hex: 75CABA4AE1BECC55FC5003 with 92 bits, messagestart 0
2017.11.04 11:35:03 4: Hideki_Parse radino_Stick_433Mhz incomming P12#75CABA4AE1BECC55FC5003
2017.11.04 11:35:03 4: Hideki_Parse SensorTyp = 30 decodedString = 755ecede23c254ff04f005
2017.11.04 11:35:03 4: radino_Stick_433Mhz decoded Hideki protocol model=Hideki_30, sensor id=5e, channel=2, bat=ok, temp=22.3, humidity=54
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:101]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:12]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:23]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:32]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:34]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:41]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:14]
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ:  +seq[:21]

>2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ: MC;P1=440;P2=-1018;P3=943;P4=-530;D=01232341412141432321434121432341412321432321432143214141434141214341414141232141434121434123232323214141434141414141214141414323214;

>2017.11.04 11:35:03 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1018;LH=943;SL=-530;SH=440;D=51D668B5B78C1732ABC0FAC;C=488;L=90;R=46;
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12 clock 488 RSSI -51 -> Hideki protocol
2017.11.04 11:35:03 4: Arduino_Nano_433Mhz: hideki protocol converted to hex: 75CABA4AE1BECC55FC5003 with 92 bits, messagestart 0

sidey79, 2017-11-05:
Hallo @HomeAutoUser 

ich mache das aktuell so, dass ich Tests in einem Unittest definiere. (Läuft nur auf einem Mega).

Dort teste ich vom Prinzip nur die Library. Es sind so ca. 30 Tests die dort duchlaufen. Wenn ich etwas an dem Code ändere, dann lasse ich alle Tests durchlaufen.
Manchmal gibt es dann gewollte Abweichungen, manchmal weiss ich dann auch nicht ob es vor oder nachher richtiger ist :)

Vom Prinzip könnte man die Lib auch auf dem PC testen. Dazu wäre eigentlich nur die Serielle Ausgabe anzupassen.
 
Alternativ hatte ich auch mal gedacht, ich könnte mit einem Arduino senden und mit dem anderen empfangen, aber das wird leider doch zu kompliziert, da ich alle Ergebnisse erst mal manuell auswerten musste.

Was die MC Ausgabe mit der Signalfolge angeht, so handelt es sich hierbei um eine Zusätzliche Debug Ausgabe. Die habe ich aktiviert, um mehr Daten zu haben, wenn ein Signal als MC erkannt wurde.

Zum Hideki Protokoll, da bräuchte ich mal mehr Daten. Erklären kann ich mir das auf Anhieb nicht.
Vor allem fehlen in dem Branch den Ralf angelegt hat noch ein paar Korrekturen. Er war da meiner Meinung nach etwas zu schnell :)

Ralf9, 2017-11-06:
Bei Testen mit meiner aktuellen Version ist mir aufgefallen, daß der Hideki-Sensor bei einer RSSI von ca 38 - 45 recht gut empfangen wird. Ab einer RSSI von ca 50 wird er nur noch sporadisch empfangen.
Wenn ich dem RXB6 als Empfänger verwende wird der Hideki-Sensor viel besser empfangen.
@HomeAutoUser
Mit welcher RSSI empfängst Du Deine Hideki-Sensoren?

Ralf9, 2017-11-06:
Hier sind die Daten die ich innerhalb einer halben Stunde empfange. Für mich ist das ausreichend.
```
2017-11-07_00:09:16 Hideki_30_b8.4 T: 22.9 H: 56
2017-11-07_00:10:02 Hideki_30_b8.4 T: 22.9 H: 55
2017-11-07_00:13:18 Hideki_30_b8.4 T: 22.8 H: 55
2017-11-07_00:13:21 Hideki_30_b8.4 T: 22.8 H: 55
2017-11-07_00:14:57 Hideki_30_b8.4 T: 22.9 H: 55
2017-11-07_00:16:34 Hideki_30_b8.4 T: 22.8 H: 55
2017-11-07_00:22:18 Hideki_30_b8.4 T: 22.8 H: 56
2017-11-07_00:23:10 Hideki_30_b8.4 T: 22.8 H: 56
2017-11-07_00:23:57 Hideki_30_b8.4 T: 22.8 H: 56
2017-11-07_00:24:48 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:26:22 Hideki_30_b8.4 T: 22.8 H: 56
2017-11-07_00:26:23 Hideki_30_b8.4 T: 22.8 H: 56
2017-11-07_00:28:01 Hideki_30_b8.4 T: 22.8 H: 56
2017-11-07_00:28:49 Hideki_30_b8.4 T: 22.8 H: 56
2017-11-07_00:29:38 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:30:27 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:31:16 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:31:17 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:32:05 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:32:09 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:33:43 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:33:43 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:34:32 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:34:32 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:35:21 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:35:25 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:36:10 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:36:59 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:36:59 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:37:03 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:37:48 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:39:27 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:41:04 Hideki_30_b8.4 T: 22.7 H: 56
2017-11-07_00:41:57 Hideki_30_b8.4 T: 22.7 H: 56
```

HomeAutoUser, 2017-11-07:
Hallo @Ralf9 | @sidey79 
ich habe es nochmal verglichen und sehe noch nicht ganz durch.
Der Empfang kippt bis zu teilweise 2/3 schlechter als die vorherigen FW bzw. diese von Ralf.
Eine genaue Erklärung finde ich noch nicht. Es ist teilweise nur bei "vereinzelten Sensoren", als ob dies mit dem RSSI zu tun hat. Beim Downgrade zur anderen FW laufen alle wieder gleich gut. *grübel*

sidey79, 2017-11-07:
@HomeAutoUser 

Verstehe ich ehrlich gesagt immer noch nicht.
Was genau ist die vorherige. Bis zu welchem commit klappt es denn besser und was genau ist mit besser gemeint?

HomeAutoUser, 2017-11-09:
Hallo, hier eine Tabelle welche das wiederspiegelt. Die genauere Einkrenzung versuche ich irgendwie mit meinen Softwareständen einzugrenzen.

[image]

sidey79, 2017-11-09:
Du kannst jeden commit auschecken und compilieren ja. :)

Mir fällt aber nicht ein, wieso die Empfangswerte etwas mit der FW zu tun haben sollten.
Diese werden ja vom cc1101 nur ausgelesen. Die ließen sich nur über andere Werte wie Empfindlichkeit beeinflussen, aber diese Werte stehen im EEPROM und nicht in der FW.

Was die sonstigen Angaben in deiner Tabelle so bedeuten müsstest Du mir noch erklären :)

Ralf9, 2017-11-10:
@HomeAutoUser 
Die ca -90 dB vom Hideki Sensor sind zu schlecht. Ich habe bei meinen Hideki Sensoren nur bei RSSI Werten besser als ca -55 dB einen guten Empfang.
Hast Du es schon mal versucht wenn Dein Hideki Sensor im Bad oder Schlafzimmer ist?
Welche Protokoll Ids haben Deine Sensoren?

Bei den Werten in Deiner Tabelle sind wahrscheinlich auch die Wiederholungen mit dabei.
Mit einem event-min-interval von ca 10 ist es wahrscheinlich aussagekräftiger, aber  das was Du damit aussagen willst, ist damit auch erkennbar.
Demnach sieht der Empfang mit meiner Version recht gut aus. Die vielen kleinen Fehlerbeseitigungen und Optimierungen scheinen sich positiv auszuwirken. Für mich passt es erstmal so.
Beim Manchester warte ich erst mal ab.
Ich werde jetzt mal einige Empfangtests machen und mich um andere Dinge kümmern.

sidey79, 2017-11-10:
@Ralf9

Wenn ich das Problem richtig verstanden habe, dann werden die Sensoren ja als MC Signal erkannt.
Du hast einige Korrekturen nicht upstream übernommen, aber am MAC Decoder hast Du doch nichts verändert oder?

Für mich wird da gerade kein Schuh draus

HomeAutoUser, 2017-11-10:
@Ralf, du hast es richtig erkannt!
Es macht den Eindruck, das der CC1101 mit einer Änderung auf einmal unempfindlicher bis "taub" gewurden ist. Ich mache mir nun die Arbeit und gehe sämtliche Commit´s durch und teste. Soeben bin ich beim Stand nach dem MC Fix im August. Hier empfange ich auch noch Sensoren mit ü90dB beginnend und regelmäßig. Das ist mit dem jetzigen Stand leider nicht so, wo auch der gesamte Empfang darunter leidet".

Ralf9, 2017-11-10:
@sidey79 
Ja, beim MC Decoder habe ich nichts geändert, durch die zusätzliche Protokoll ID 12.1 funktioniert er bei den Hideki recht gut. Bei etwas schlechterm Empfang sind die MC-Nachrichten zu kurz. Wenn ich das MC deaktiviere sind auch die MU-Nachrichten des Hideki zu kurz.

@HomeAutoUser 
Sensoren mit den Protokoll Ids 0 und 7 empfange ich auch bei einer RSSI mit ca -85 bis -92 dB regelmässig.
```
2017.11.10 22:41:35.176 3 : sduino Dispatch: P7#5A10D3F37, -91.5 dB, dispatch
2017-11-10 22:41:35.180 SD_WS07 SD_WS07_TH_5A2 T: 21.1 H: 55

2017.11.10 22:46:17.584 3 : sduino Dispatch: s7190BA79A800, -92.5 dB, dispatch
2017-11-10 22:46:17.585 CUL_TCM97001 GT_WT02_88 T: 18.6 H: 60
```
Ich habe den Eindruck, daß der cc1101 empfindlicher gegenüber Störungen ist als ein RXB6.
Ich habe mal den cc1101 sduino per wlan abgesetzt.
Hier sind die empfangenen Nachrichten bei einer RSSI von -66 dB

```
2017-11-10_22:22:16 Hideki_30_b8.4 T: 19 H: 56
2017-11-10_22:22:16 Hideki_30_b8.4 T: 19 H: 56
2017-11-10_22:24:43 Hideki_30_b8.4 T: 18.7 H: 56
2017-11-10_22:26:21 Hideki_30_b8.4 T: 18.5 H: 56
2017-11-10_22:26:21 Hideki_30_b8.4 T: 18.5 H: 56
2017-11-10_22:27:10 Hideki_30_b8.4 T: 18.5 H: 56
2017-11-10_22:27:59 Hideki_30_b8.4 T: 18.4 H: 56
2017-11-10_22:28:48 Hideki_30_b8.4 T: 18.4 H: 56
2017-11-10_22:29:37 Hideki_30_b8.4 T: 18.3 H: 56
2017-11-10_22:30:26 Hideki_30_b8.4 T: 18.3 H: 56
2017-11-10_22:31:15 Hideki_30_b8.4 T: 18.2 H: 56
2017-11-10_22:31:15 Hideki_30_b8.4 T: 18.2 H: 56
2017-11-10_22:32:04 Hideki_30_b8.4 T: 18.2 H: 56
2017-11-10_22:32:53 Hideki_30_b8.4 T: 18.1 H: 56
2017-11-10_22:32:53 Hideki_30_b8.4 T: 18.1 H: 56
2017-11-10_22:36:09 Hideki_30_b8.4 T: 18 H: 57
2017-11-10_22:36:58 Hideki_30_b8.4 T: 17.9 H: 57
2017-11-10_22:37:47 Hideki_30_b8.4 T: 17.9 H: 57
2017-11-10_22:37:47 Hideki_30_b8.4 T: 17.9 H: 57
```

sidey79, 2017-11-11:
Also wenn Du nichts verändert hast, woran soll es dann liegen? Der MC Decoder hat Priorität vor den MU Ausgaben.

Übernimmst Du die Korrekturen aus dem Hauptprogramm Repo noch?
Den Zweck deines Forks habe ich auch noch nicht so ganz verstanden.

HomeAutoUser, 2017-11-11:
Moin Moin,
nach allen Testmöglichkeiten, bin ich nun zu dem Entschluss gekommen, d**as es am besten funktioniert vor dem fix MC**, FW [diese](https://github.com/RFD-FHEM/SIGNALDuino/tree/1f56bbf9f300437567a78aecf457700f66a08aac). Da werden die Hidekiprotokolle je Sendung bis zu 3x erkannt, da diese auch mehrfach senden. **Das ganze ist ein MC - Protokoll.** Wenn wir danach bisher nur 1 Wiederholung erkennen und somit das ggf. eine Verschlechterung darstellt ist die kein Schritt vor was den MC Decoder darstellt. Das der Empfänger anders arbeitet als ein LowBudget, klar, aber es zeigt die Anwendung, wenn ich die andere FW nutze, so klappt es besser. Was genau geändert wurde, da kann ich natürlich nicht mitreden, da ich im Detail nicht darin vertieft bin.

Der Zweck diesen Fadens, dient dazu, klar zu machen, wenn ein Sensor bis zu 3 Wiederholungen sendet, das es auch Ziel sein sollte, dieses Soll zu erreichen. Ja, das SOLL ist eine theroretische Angabe welche wir aus Zeit und Sendevorgaben berechnen können, dennoch sollten wir uns daran orientieren. Wenn ich mit einer anderen FW ohne Veränderung der Position bzw. HW, ein besseres Verhältnis zwischen SOLL und IST Werten bekomme, so sollte dies auch wiederspiegeln, das es am Ablauf liegen sollte.

Sollten wir nun den "Weg" einschlagen, das nach hinten zu kehren weil ein LowBudget besser funktioniert, so schlagen wir erneut den Weg ein, eine spezifische FW zu "bauen".

>Ich habe den Eindruck, daß der cc1101 empfindlicher gegenüber Störungen ist als ein RXB6.

Der CC1101 ist empfindlicher, deswegen empfange ich auch mehr über diesen aber als Störungen sehr ich dies nicht.

elektron-bbs, 2017-11-11:
Diese Firmware hatte aber den Haken, das die MC-Erkennung komplett abgeschaltet werden musste, um z.B. Sensoren der WS2000-Serie zu empfangen. Verschiedene Empfänger zu vergleichen bringt auch wenig, da der eine in einer Empfangssituation besser ist und der andere vielleicht woanders. Habt ihr euch eigentlich schon mal die Mühe gemacht und mit den Einstellungen des CC110x zu experimentieren? 
Ich habe da mal irgendwann verschiedene Tests gefahren. Hier z.B. der Einfluss der Bandbreite auf den Empfang:
[image]
Eine Verringerung der Bandbreite bringt prinzipiell erst mal besseren Empfang, weil der Empfänger weniger Störungen einfängt. Allerdings wird der Empfang von Sensoren, die nicht genau auf der Sollfrequenz senden dann schlechter.
Interessant auch die Einstellung des sens-Wertes auf die gemessenen Low-/High-Zeiten:
[image]
Gerade diese Einstellung könnte für Manchester ausschlaggebend sein, weil der Takt aus diesen Zeiten ermittelt wird.

sidey79, 2017-11-11:
@HomeAutoUser

Bin noch unsicher, wie sich deine Erkenntnisse zusammen setzen.

In der Tabelle hast Du verschiedene FW Stände verglichen. 
Vermutlich hast Du auch verschiedene uC und Empfänger verwendet.

Bei den ersten beiden Tests in deiner Tabelle hast Du eine FW mit angepasstem MC Decoder angegeben.

Bei dem 3. Test hast Du kein commit angegeben, allerdings nehme ich an, dass auch dort der MC Fix bereits enthalten ist.

An den RSSI Werten ändert der MC Decoder nichts. Der Wert kommt vom cc1101 und kann nur über die Register beeinflusst werden. Es ist auch eher eine Information für den Anwender, der Code verwertet diese Information nicht.

Ich frage mich wie wir hier weiter kommen, der MC Decoder war früher fehlerhaft. Das habe ich angepasst.
Leider tippe ich gerade im Dunkeln, was dazu führt, dass weniger erkannt wird.
Unklar ist mir leider völlig, was mit den Daten passiert, die nicht als Hideki erkannt werden.
Werden diese als MC oder als MU übergeben oder überhaupt nicht?

Gibt es zwischen den Wiederholungen eine Sendepause oder hängen diese direkt aneinander?

Ralf9, 2017-11-11:
evtl macht es auch Sinn die Nachrichten vom Hideki Sensor mit deaktiviertem MC Decoder anzuschauen, wenn Sie schon als MU-Nachrichten unbrauchbar (z.B. zu kurz) sind, dann kann der MC decoder auch nichts sinnvolles mehr draus machen.

So sehen die Hideki als MU-Nachrichten aus, wenn sie gut empfangen werden:
```
MU;P0=-32001;P1=926;P2=-1024;P3=-528;P4=444;D=0121213434243434343121342434312134342124312124313421343434342434343431343434342124343431243134212121212434343124312434313424343124343;CP=4;R=55;
MU;P0=-6912;P1=936;P2=-1021;P3=-533;P4=446;D=0121213434243434343121342434312134342124312124343121343434342434343431343434342124343431243134212121212434343124313424312431342434343;CP=4;R=56;
MU;P0=-32001;P1=940;P2=-1032;P3=-541;P4=435;D=0121213434243434343121342434312134342124312124312431343434342434343431343434342124343431243134212121212434343124312121343434243434343;CP=4;R=55;
```

elektron-bbs, 2017-11-12:
@sidey79 

> Gibt es zwischen den Wiederholungen eine Sendepause oder hängen diese direkt aneinander?

Crestaprotocol.pdf:
"All sensors seem to repeat the same package 3 times during each data transfer. The packages are separated by a pause of at least 10mS but no longer than 50mS."

Nach welchen Kriterien trennt der SIGNALduino denn jetzt die Nachrichten?
Bei MU habe ich bei der aktuellen Firmware beobachtet, das er FS20 meistens in 3 Wiederholungen teilt. Bei FHT macht er es nicht. Dort ist es aber immer nur eine Wiederholung. Die Pause bei beiden Protokollen beträgt jeweils ca. 10 mSek.

sidey79, 2017-11-12:
Halten sich die Hersteller der Sensoren auch an diese Beschreibung des Cresta Protokolls?

Die 10 MS Pause zwischen den Übertragungen dürften dazu führen, dass jede Wiederholungen eigenständig erkannt wird. Eventuell ist die 1. und 3. nicht vollständig. Die 2. sollte es wohl sein.

Zu den MU Nachrichten:
Diese werden nicht getrennt, da es kein Merkmal gibt, wodurch diese getrennt werden.
Entweder der Musterpuffer ist voll oder der Nachrichtenpuffer oder es wird eine lange Sendepause erkannt. Dann wird die Nachricht ausgegeben.

elektron-bbs, 2017-11-12:
Tja, und wie lang muss denn nun die Sendepause sein, damit er die Wiederholungen einzeln ausgibt?

Hier mal zwei Beispiele, das die FS20-Nachrichten getrennt werden:
[Log.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/1464602/Log.txt)

und ein Auszug aus dem Log vom FHT, wobei die Nachricht nicht getrennt wird:
[Log_FHT.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/1464612/Log_FHT.txt)

Was mich auch wundert, das neuerdings manchmal die Startzeit und die Pausenzeit (P0 und P5) mit ausgegeben wird und manchmal nicht:
MU;P0=-32001;P1=383;P2=-410;P3=572;P4=-619;P5=-9288;D=01212121212121212121212123412...
MU;P1=392;P2=-392;P3=589;P4=-599;D=01212121212121212121212123412341234121234123412...

sidey79, 2017-11-12:
Bei der 2. Nachricht würde ich tippen, dass der Nachrichten Puffer voll war und ein O am Ende steht.

elektron-bbs, 2017-11-12:
Das ist richtig, aber die Frage war: 

> Tja, und wie lang muss denn nun die Sendepause sein, damit er die Wiederholungen einzeln ausgibt?

Bei FS20 wird insgesamt 3 mal mit jeweils einer Pause von 10 mSek gesendet. Da lief bisher natürlich immer der Puffer voll. Seit dem letzten Update werden diese Nachrichten aber geteilt ausgegeben und es erfolgt kein Overflow. Mir scheint, als ob einen Teil der Auswertung schon der MC-Dekoder übernimmt und zumindest schon die Trennung übernimmt.

sidey79, 2017-11-12:
Damit die Ausgabe getrennt erfolgt muss die Pause 30 ms oder länger andauern.

Ein Auftrennen von Wiederholungen bei einem Signal ohne Sync wird nicht möglich sein. So etwas ist in der aktuellen Firmware auch nicht enthalten.

Aktuell geht das nur bei Nachrichten mit einem Sync. Der Sync setzt sich aus 1*Clock und -n*Clock zusammen. Daran wird der Beginn einer Nachricht erkannt.

Was man machen könnte ist

closed 2017-11-12 by sidey79

HomeAutoUser, 2017-11-12:
Guten Abend,
da ich am WE nicht dazu kam, nun mal noch ein paar Worte von mir.
>Vermutlich hast Du auch verschiedene uC und Empfänger verwendet.

Da liegst du zum teil verkehrt. Die Tabelle wurde aufgezeichnet von deinem radino mit cc1101 als "Vergleich" um zu sehen was überhaupt kommt. Sämtliche Gliederungen die untereinander aufgelistet sind via den FW Ständen sind ALLE vom gleichen EMPFÄNGER. Deswegen kam ich ja darauf, das dort ein Wurm ist. Es kann nicht sein, bei gleicher Technik, das es auf einmal so und mal so geht mit änderung der FW.

>Unklar ist mir leider völlig, was mit den Daten passiert, die nicht als Hideki erkannt werden.
Werden diese als MC oder als MU übergeben oder überhaupt nicht?

Ich werde diesbezüglich nun mal einen Test fahren. Ich werde vorerst alles als MU aufzeichnen und dann mir die Mühe machen und schauen wie das aussieht vom Empfang her. Das selbige werde ich mit einem anderen FW stand machen. Wenn ich erkennen sollte, das der Zeitraum der MU nachrichten gleich ist, so lässt sich vermuten, das diese nicht als MC erkannt werden.

HomeAutoUser, 2017-11-12:
Moin Moin, mir fiel gerade etwas auf.
@sidey79 und @Ralf9, schaut Euch bitte mal diese Meldungen an.

Einmal werden diese dekodiert weil der P1 positiv ist und bei der selben Nachricht auf einem anderen Empfänger ist P1 negativ und wird nicht verarbeitet. Es handelt sich nun hier um ein und die selbe FW. (aktuelle dev-Version). Das Verhalten ist nicht Empfänger gebunden. Es gibt auch Fälle, da empfängt der eine mal nix und der andere nicht. Ist das so korrekt???

>2017.11.13 00:09:07 4: Arduino_Nano_433Mhz/msg READ: MC;P1=942;P2=-1013;P3=-526;P4=459;D=0121213434243121212431243121343421243121243134243434343434343434343134343434212134342134243431212121243124313434243434343434313424343;
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1013;LH=942;SL=-526;SH=459;D=51AB68B59FFC144EAB63F9C;C=489;L=90;R=38;
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz: Found manchester Protocol id 10 clock 489 RSSI -55 -> OSV2o3
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12 clock 489 RSSI -55 -> Hideki protocol
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz: hideki protocol converted to hex: 7595BACA00BE3755393003 with 92 bits, messagestart 0
2017.11.13 00:09:07 4: Hideki_Parse Arduino_Nano_433Mhz incomming P12#7595BACA00BE3755393003
2017.11.13 00:09:07 4: Hideki_Parse SensorTyp = 30 decodedString = 75bfce5e00c259ff4b5005
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz decoded Hideki protocol model=Hideki_30, sensor id=bf, channel=4, bat=ok, temp=20, humidity=59
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12.1 clock 489 RSSI -55 -> Hideki protocol not invert
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz: Found manchester Protocol id 52 clock 489 RSSI -55 -> OS_PIR
2017.11.13 00:09:07 4: Arduino_Nano_433Mhz: Found manchester Protocol id 58 clock 489 RSSI -55 -> tfa 30.3208.0 

>2017.11.13 00:09:07 4: radino_Stick_433Mhz/msg READ: MC;P1=-1005;P2=936;P3=-525;P4=451;D=012123434143212121432143212343412143212143234143434343434343434343234343434121234341234143432121212143214323434143434343434323414343;
2017.11.13 00:09:07 4: radino_Stick_433Mhz/msg READ: MC;LL=-1005;LH=936;SL=-525;SH=451;D=A356D16B3FF8289D56C7F38;C=486;L=89;R=38;
2017.11.13 00:09:07 4: radino_Stick_433Mhz: Found manchester Protocol id 10 clock 486 RSSI -55 -> OSV2o3
2017.11.13 00:09:07 4: radino_Stick_433Mhz: Found manchester Protocol id 12 clock 486 RSSI -55 -> Hideki protocol
2017.11.13 00:09:07 4: radino_Stick_433Mhz: Found manchester Protocol id 12.1 clock 486 RSSI -55 -> Hideki protocol not invert
2017.11.13 00:09:07 4: radino_Stick_433Mhz: Found manchester Protocol id 58 clock 486 RSSI -55 -> tfa 30.3208.0 

>2017.11.13 00:12:39 4: Arduino_Nano_433Mhz/msg READ: MC;P1=930;P2=-1027;P3=-532;P4=438;D=012121343424343121243134243121343421243121243134212134243434343431343434342121342431342434312121212431343434213424313434243134243;
2017.11.13 00:12:39 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-1027;LH=930;SL=-532;SH=438;D=28EB345ACA7E0A675584C66;C=487;L=91;R=42;
2017.11.13 00:12:39 4: Arduino_Nano_433Mhz: Found manchester Protocol id 10 clock 487 RSSI -53 -> OSV2o3
2017.11.13 00:12:39 4: Arduino_Nano_433Mhz: Found manchester Protocol id 12 clock 487 RSSI -53 -> Hideki protocol
2017.11.13 00:12:39 4: Arduino_Nano_433Mhz: hideki protocol converted to hex: 75CABACA0DBE33556F6701 with 91 bits, messagestart 1
2017.11.13 00:12:39 4: Hideki_Parse Arduino_Nano_433Mhz incomming P12#75CABACA0DBE33556F6701
2017.11.13 00:12:39 4: Hideki_Parse SensorTyp = 30 decodedString = 755ece5e17c255ffb1a903
2017.11.13 00:12:39 4: Arduino_Nano_433Mhz decoded Hideki protocol model=Hideki_30, sensor id=5e, channel=2, bat=ok, temp=21.7, humidity=55

>2017.11.13 00:12:39 4: radino_Stick_433Mhz/msg READ: MC;P1=-1015;P2=932;P3=-515;P4=455;D=01212343414343212143234143212343412143212143234121234143434343432343434341212341432341434321212121432343434123414323434143234143;
2017.11.13 00:12:39 4: radino_Stick_433Mhz/msg READ: MC;LL=-1015;LH=932;SL=-515;SH=455;D=A3ACD16B29F8299D5613198;C=486;L=89;R=45;
2017.11.13 00:12:39 4: radino_Stick_433Mhz: Found manchester Protocol id 10 clock 486 RSSI -51.5 -> OSV2o3
2017.11.13 00:12:39 4: radino_Stick_433Mhz: Found manchester Protocol id 12 clock 486 RSSI -51.5 -> Hideki protocol
2017.11.13 00:12:39 4: radino_Stick_433Mhz: Found manchester Protocol id 12.1 clock 486 RSSI -51.5 -> Hideki protocol not invert
2017.11.13 00:12:39 4: radino_Stick_433Mhz: Found manchester Protocol id 58 clock 486 RSSI -51.5 -> tfa 30.3208.0 
2017.11.13 00:12:39 4: radino_Stick_433Mhz/msg READ: MC;P1=942;P2=-525;P3=456;P4=-1011;D=0123432323232321232323234141234321234323214141414;

elektron-bbs, 2017-11-13:
@sidey79 

> Ich habe auch mit einer Baudrate von 500Kbaud getestet. Das bringt doch einen erheblichen Booster, ob das bei allen so läuft, weiss ich nicht.

Wahrscheinlich nicht. Ich habe es gerade 250k und 500k auf einem Pro Mini getestet: Die Schnittstelle geht nach kurzer Zeit wieder auf "closed". Dann habe ich mir die Doku von dem verwendeten PL2303 angesehen, dort sind diese Baudraten auch nicht vorgesehen:
[image]
Beim CH340 sieht es auch nicht besser aus:
"CH340 supports common baud rates: 50, 75,
100, 110, 134.5, 150, 300, 600, 900, 1200, 1800, 2400, 3600, 4800, 9600, 14400, 19200,
28800, 33600, 38400, 56000, 57600, 76800, 115200, 128000, 153600, 230400, 460800,
921600, 1500000, 2000000 baud."

sidey79, 2017-11-13:
Der CH340 rennt bei mir seit tagen :)

sidey79, 2017-11-13:
@elektron-bbs 

Du kannst ja mal den enhanced-processtime Branch probieren.

sidey79, 2017-11-14:
@elektron-bbs 

Wo kommt der PL2302 überhaupt zum Einsatz? Der ist doch weder auf einem NANO noch Pro Mini verbaut.

Grüße Sidey

elektron-bbs, 2017-11-14:
Ich habe hier einen Mini Pro mit dem PL2303 - sicher irgendein China-Clone. In meiner Sammlung befindet sich auch noch ein Nano mit einem gefälschten FT232. Man ist da vor nichts sicher.
Ich teste jetzt gerade den enhanced-processtime Branch. Der Mini Pro 3,3V 8MHz mit einem CC1100 empfängt damit gar nichts. Mit dem Radino funktioniert die Firmware.

sidey79, 2017-11-14:
Mini Pro? Meinst Du Pro Micro oder Pro Mini?

Der Pro mini sollte so etwas ja nicht haben.

elektron-bbs, 2017-11-14:
Sorry, ich habe da Blödsinn geschrieben. Es handelt sich um einen Pro Mini - natürlich ohne USB. Der PL2303 sitzt auf einem extra Adapter.

HomeAutoUser, 2017-11-14:
@sidey79,
zwecks der Hidekierkennung, ist das hier vielleicht ein Weg um ggf. dort Erkenntnisse zu finden?
Ich habe mir mal die Mühe gemacht alle Daten herauszufiltern welche einen Overflow erzeugen und somit vermutlich auch die Protokolle hintereinandner übertragen werden. Vermutlich ist ja in der Trennung / Erkennung noch Feinschliff angebracht.

>2017.11.14 23:15:57 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-2788;P1=924;P2=-1048;P4=-525;P5=423;D=01212121212121212121212121212121214525414525414521212541452541452541452541212121214521254145254145254121452125414525412145254145212125412145254145254145254145254145254145212125412145212125414521212125414525414521212541214525414525414521254121214525414521;CP=5;R=219;O;
2017.11.14 23:15:57 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-1046;P1=112;P2=-9924;P3=75;P5=-530;P6=446;D=01230303030303030303030303035603065356065303030356012303030303030303030303030303030303560653560653560303065356065356065356065303030303560306535606535606530356030653560653035606535603065303035606535606535606535606535606535603030653035603030653560303030653;CP=3;R=219;O;
2017.11.14 23:17:08 4: Arduino_Nano_433Mhz/msg READ: MU;P0=712;P1=932;P2=-1026;P4=-520;P5=432;P6=116;P7=-2228;D=12121212121212121212121212121212145254145254121452541214521254121452125412121214521254145254145254121452125414525412145212121212541452121254121452121212121212121254121452125412121452121254121212145254121452541452541212121452541452126702121212121212121212;CP=1;R=218;O;
2017.11.14 23:18:19 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-32001;P1=931;P2=-1025;P4=-535;P5=422;P6=-2308;P7=720;D=01212121212121212121212121212121214525414525414521254121452125412145212541212121452125414525414525412145212541452541214521212121254145212125412145212121212121212125412145212541212145212125412145254145254121452541452121212125412121216721212121212121212121;CP=1;R=219;O;
2017.11.14 23:19:30 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-19044;P1=179;P2=-1035;P4=-534;P5=414;P6=-2388;P7=708;D=01212121212121212121212121212121214525414525412121452541452125412145212541212121452125414525414525412145212541452541214521212121254145212125412145212121212121212125412145212541212145212125412121452125412145254145254121452121254121216721212121212121212121;CP=1;R=219;O;
2017.11.14 23:21:52 4: Arduino_Nano_433Mhz/msg READ: MU;P0=684;P1=916;P2=-1035;P4=-531;P5=408;P6=112;P7=-2288;D=12121212121212121212121212121212145254145254121452125414521254121452125412121214521254145254145254121452125414525412145212121212541452121254121452121212121212121254121452125412121452121254121212121452541452541452125414525412145254145267021212121212121212;CP=1;R=219;O;
2017.11.14 23:23:03 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-4356;P1=918;P2=-1048;P4=-535;P5=426;D=01212121212121212121212121212121214525414525414521212541452541452541452541212121214521254145254145254121452125414525412145254145212125414521212121212541214521212541452125412145212125414521212125414525414521212541214525414525414521254121214525414521212541;CP=5;R=218;O;
2017.11.14 23:23:03 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-1059;P1=903;P2=-533;P3=429;P4=112;P5=-9972;P6=696;D=01010101010101010101010101210101230101032123045601010101010101010101010101010101230321230321230101032123032123032123032101010101230103212303212303210123010321230321012303212301032101230101010101032101230101032123010321012301010321230101010321230321230101;CP=1;R=219;O;
2017.11.14 23:24:14 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-1031;P1=421;P2=-530;P3=917;P4=92;P5=-2160;P6=684;D=01230303030303030303030303030303032101232101230321012303210301230321030123030303210301232101232101230321030123210123032103030303012321030301230321030303030303030301230321030123030321030301230303032101230321012321012303030321012321030456030303030303030303;CP=3;R=218;O;
2017.11.14 23:26:36 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-20692;P1=918;P2=-1031;P3=-525;P4=423;P5=-2444;P6=708;D=01212121212121212121212121212121213424313424312121342431342124312134212431212121342124313424313424312134212431342431213421212121243134212124312134212121212121212124312134212431212134212124312121342124312134243134243121342121243121215621212121212121212121;CP=1;R=220;O;
2017.11.14 23:28:12 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-1002;P1=472;P2=-1942;P3=-3904;P4=116;P5=-16476;P6=956;P7=-509;D=01010101010101010121210101210121212121010101012101210131210101010101010101010101010101010101212101012101212121210101010121012104560606717101767171717106710176067171060176060176710606717171710171767171717106017171717171767106017176060176067171717171060671;CP=1;R=232;O;
2017.11.14 23:30:09 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-6184;P1=935;P2=-1041;P4=-527;P5=423;D=01212121212121212121212121212121214525414525414521212541452541452541452541212121214521254145254145254121452125414525412145254145212121212541452121212121212125412145212541214521212541452121212541452541452121254121452541452541452125412121452541452121254121;CP=5;R=219;O;
2017.11.14 23:30:09 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-1076;P1=878;P2=-537;P3=415;P4=116;P5=-9888;D=01010101010101010101010101010121010123032123045101010101010101010101010101010101230321230321230101032123032123032123032101010101230103212303212303210123010321230321012303212301032123010321230101010101010103210123010321012301010321230101010321230321230101;CP=1;R=218;O;
2017.11.14 23:31:20 4: Arduino_Nano_433Mhz/msg READ: MU;P0=668;P1=910;P2=-1057;P4=-527;P5=432;P6=92;P7=-2260;D=1212121212121212121212121212121214525414525412145254121452125412145212541212121452125414525414 [... cut ...] 145212121212121212125412145212541212145212125412121452125412145254145212541452541452125412167212121212121212121;CP=1;R=3;O;
2017.11.14 23:33:42 4: Arduino_Nano_433Mhz/msg READ: MU;P0=419;P1=-1053;P2=926;P3=-538;P4=-2416;P5=720;D=01212121210103212301212121032301212121032123012121212121212121032123012103212123012121032123012121032123010323010323012103212301032124512121212121212121212121212121212301032301032121230103230121032123012103212121230121032301032301032123012103230103212301;CP=0;R=217;O;

sidey79, 2017-11-15:
Danke, die Daten sehen gut aus. Bei vielen sieht man auch, den Beginn einer weiteren Übertragung, ich denke das hilft.

sidey79, 2017-11-15:
@HomeAutoUser 
Ich war vielleicht etwas zu voreilig. Die Daten scheinen alle eher dem Oregon V2 Format zu entsprechen.
Das erkannt man an der Abfolge der langen Pulse -1000, +900.

Hat dein Hideki Sensor vielleicht eine Taste die Du drücken kannst?

HomeAutoUser, 2017-11-15:
@sidey79 

>Hat dein Hideki Sensor vielleicht eine Taste die Du drücken kannst?

Sie verfügen maximal über eine "MiniReset-Taster".

 >Ich war vielleicht etwas zu voreilig. Die Daten scheinen alle eher dem Oregon V2 Format zu entsprechen. Das erkannt man an der Abfolge der langen Pulse -1000, +900.

Sollte ich falsch geschaut haben, so werde ich gern nochmal die Zeiten vergleichen und auch ggf, gegenprüfen mit einer "nur MU" Protokollaufnahme.

Ralf9, 2017-11-15:
@HomeAutoUser
Ist dies Dein Oregon V2 Sensor oder einer vom Nachbar?
Hattest Du den MC-Decoder ausgeschaltet?
Wenn ich den Code in der ManchesterpatternDecoder::doDecode() richtig verstehe, dann müsste eigentlich das Nachrichtenende erkannt und auch die Wiederholungen ausgegeben werden.

Ralf9, 2017-12-24:
Ich wünsche Euch allen frohe Weihnachten

sidey79, 2017-12-24:
Auch von mir ein frohes Fest.