RFD-FHEM/SIGNALDuino #69: Erweiterung für 2. Conrad RSL Modul

litronics, 2017-11-04:
Ich habe das aktuelle RSL Modul so angepasst, dass es mit den Conrad RSL Schaltern funktioniert und als eigenes Modul implementiert.

Für das Modul gibt es auch schon einen Thread im Forum:
[https://forum.fhem.de/index.php/topic,78904.0.html](https://forum.fhem.de/index.php/topic,78904.0.html)

Damit das funktioniert benötige ich allerdings drei Anpassungen in dem SIGNALduino Modul.

Folgende Variablen müssten zusätzliche Einträge erhalten:

%matchListSIGNALduino -->
`"75:SD_RSL2" => "^P1RSL2#[A-Fa-f0-9]{8}",`

$clientsSIGNALduino  --> 
`."SD_RSL2:"`

%ProtocolListSIGNALduino  -->
`"75" => 
   {
      name => 'ConradRSL2', 
      id => '75',
      one => [3,-1],
      zero => [1,-3],
      clockabs => 500, 
      format => 'twostate', 
      preamble => 'P1RSL2#', 
      clientmodule => 'SD_RSL2', 
      modulematch => '^P1RSL2#[A-Fa-f0-9]{8}', 
      length_min => '32',
   }, `

sidey79, 2017-11-05:
Danke für den Beitrag.

Vielleicht lässt sich das ja auch in das vorhandenen RSL Modul mit einbauen

@Ralf9 was meinst Du?

litronics, 2017-11-06:
Das lässt sich theoretisch sicher machen - die zwei Module sind ja in großen Teilen identisch.

Ralf9, 2017-11-06:
Ich würde es auch besser finden, wenn man es in das vorhandene 14_SD_RSL Modul integrieren könnte.
Ich habe die Vermutung/Verdacht, daß das Conrad RSL beim EZ-Control nicht ganz sauber impementiert ist und deshalb dafür die neue Protokoll Id 75 benötigt wird.
Beim Senden mit dem SD_RSL2 Modul funktioniert es ja auch mit der Protokoll Id 1.

Sind die aktuellen Conrad RSL Schalter nicht mehr Intertechno kompatibel?
Wenn sie sich mit den ITv3 Protokoll anlernen lassen, dann könnte man das IT-Modul verwenden.

litronics, 2017-11-07:
Nope sind sie leider nicht - ich hab von IT beide Varianten versucht - aber mit keiner lassen sich die Schalter anlernen.

Ralf9, 2017-11-16:
Ich habe die neue Protokoll Id 75 eingebaut
https://github.com/RFD-FHEM/RFFHEM/commit/a26ac697b13c10a19e14896e8bbc9b93fc60c3a4

Ich habe auch im SD_RSL Modul was angepasst, dabei ist mir aufgefallen, daß die my @RSLCodes und die my @RSL2Codes fast gleich sind, sie sind nur um 2 Zeilen verschoben.

litronics, 2017-11-16:
ja, die sind super ähnlich - der Unterschied liegt nur in den ersten Bits - denn bei meiner Variante wird mit dem 5. Bit ON und OFF definiert - in der ursprünglichen ist das anders codiert.

elektron-bbs, 2018-10-14:
Ich weiß jetzt nicht, wer für das Modul zuständig ist, aber man tappt bei diesem Log-Eintrag völlig im Dunkeln, was ihn verursacht:

2018.10.14 17:22:22 3: UNDEFINED Remotebutton send to define: 111DB1_0_0

Verursacher ist wohl die 14_SD_RSL.pm - Bitte mal die Logausgabe

      Log3 $hash, 3, "UNDEFINED Remotebutton send to define: $deviceCode";

ändern in z.B.:

      Log3 $hash, 3, "$name RSL_Parse UNDEFINED Remotebutton send to define: $deviceCode";

litronics, 2018-10-16:
Ich weis nicht, was Du vor hast oder machst - aber es sieht so aus als hättest du ein Gerät mit einem Taster der bei Betätigung von dem Modul erkannt wird.

elektron-bbs, 2018-10-16:
Ich wollte nur darauf hinweisen, das man aus der Log-Ausgabe nicht erkennt, um welches Protokoll es sich handeln könnte.

HomeAutoUser, 2018-11-19:
@litronics 
bitte erfasse von deiner Remote mal mehr RAWMSG. Du hast keinen Start und so treffen auch andere Protokolldefinitionen zu. Gern würde ich der Sache noch einmal auf den Grund gehen und dort ist bestimmt Potienzial ;-)

litronics, 2018-11-20:
@HomeAutoUser
da muss ich leider passen - ich habe die Conrad RSL2 Aktoren mittlerweile ausgetauscht gegen HMIP Aktoren. Die RSL2 Aktoren waren nicht wirklich zuverlässig und mussten öfter mehrfach geschalten werden bevor sie überhaupt irgendwas gemacht haben. Mit den HMIP Aktoren waren diese Probleme sofort weg.

HomeAutoUser, 2018-11-20:
Danke für die Antwort. Da kann ich nur weiter suchen! @Ralf9 hast du solche Aktoren?

Ralf9, 2018-11-20:
Nein, ich habe kein RSL. 
So viel Potienzial dürfte die ID 75 nicht haben. Evtl macht es Sinn die ID 75 mit der developId p oder y zu deaktivieren.
Nachdem was ich in Errinnerung habe und im fhem forum gefunden habe hat @litronics keine orginal RSL Fernbedienung 
> Ich habe aber noch ein paar RSL Schalter die ich von meiner EZ-Control steuern kann.

gandi1791, 2019-01-02:
Ich denke zwar, dass sich das hier erübrigt, da ja im fhem Forum Sidey und Ralf9 auch im dem [Fred ](https://forum.fhem.de/index.php/topic,58397.msg878288.html#msg878288 )aktiv sind, in dem ich meine Probleme versuche (mit Eurer Hilfe) zu lösen.

Nur nochmal der Vollständigkeit halber.
Ich habe 5x [Funkschalter](https://www.conrad.de/de/rsl-funk-schalter-unterputz-1-kanal-schaltleistung-max-3500-w-reichweite-max-im-freifeld-70-m-1181990.html) und 1x [Fernbedienung](https://www.conrad.de/de/rsl-funk-fernbedienung-12-kanal-reichweite-max-im-freifeld-30-m-526925.html) von Conrad.