RFD-FHEM/SIGNALDuino #72: MU-Nachrichten werden z.T. als MC erkannt, z.B. Siro und andere

HomeAutoUser, 2017-11-06:
Für die Übersicht, lieber ein neuer Faden.
Immer noch sporadische fälschliche Erkennung nach https://github.com/RFD-FHEM/SIGNALDuino/commit/cfb3738f3139fa28f80314c71eb5cf4aa26896ec Anpassung (signalDecoder.cpp).

Zum Teil FHTTK oder FHT80b. da besteht noch Potential ;-)

>2017.11.06 18:21:17 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 60 dmsg K11222250 length 32 RSSI = -55
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=396;P2=-9756;P3=-393;P4=607;P5=-610;D=012131313451345451345131313451345454513131313131313131313451345131345451313131313131313131313454513454545454545;CP=1;SP=2;R=5;m=2;
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:13]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:31]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:51]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ:  +seq[:10]
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ: MC;P1=396;P3=-393;P4=607;P5=-610;D=10;
2017.11.06 18:21:30 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-610;LH=607;SL=-393;SH=396;D=FFFBDADED5FFDBAFFF5AAA8;C=334;L=89;R=5;
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz/msg READ:  +seq[:43]
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=377;P2=-412;P3=584;P4=-607;D=012121212121212121212121234341212121212123412123412121234341234121212123434341234121212121212123434343434121212341212;CP=1;R=3;
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz: FHT80TF - door/window switch protolength 45
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 70 dmsg T81460E01 length 32 RSSI = -72.5
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.06 18:22:04 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate

>2017.11.06 18:20:30 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:10]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:103]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:54]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:41]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ:  +seq[:16]
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ: MC;P1=586;P4=-406;P5=383;P6=-613;D=5;
2017.11.06 18:20:31 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-613;LH=586;SL=-406;SH=383;D=576FDD55D8;C=331;L=37;R=22;
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READ:  +seq[:41]
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READ:  +seq[:15]
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=391;P2=-400;P3=610;P4=-614;D=0121212121212121212121212341212341212121212341212341234341212341212121212121212123412341212343412121212341212343412341212341212341212121512121212121212121212121234121234121212121234121234123434121234121212121212121212341234121234341212121234121234341234;CP=1;R=33;O;
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz: FHT80 - roomthermostat protolength 54
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 73 dmsg 810c04xx0909a001202C0000A626 length 48 RSSI = -57.5
2017.11.06 18:20:42 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate

>2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=414;P3=-608;P5=-413;P7=591;D=01023434315151515151573157315157315737315157373731573737315151515731545731515151573734573737373;CP=1;R=67;
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:13]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:43]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:35]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:56]
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ: MC;P3=-607;P4=612;P5=384;P6=-402;D=5;
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-607;LH=612;SL=-402;SH=384;D=55EBAAF77B5B8;C=334;L=49;R=8;
2017.11.06 18:18:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
[... 47 lines cut ...]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:52]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:43]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:62]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ: MC;P2=-412;P3=617;P4=-605;P5=386;D=5;
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-605;LH=617;SL=-412;SH=386;D=DFDADDD755EEF7B5;C=336;L=64;R=9;
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:32]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:24]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=400;P2=-431;P4=574;P5=-603;D=012121212321212321212121245123212451245454512124512121212121245124545124512124512124545451245454532121212451212451212124532451245124512;CP=1;R=62;
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.06 18:05:17 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate

...

sidey79, 2017-11-07:
Ich habe mal was versucht: 4ed6c3b

Problem ist vermutlich, dass ich den Takt erst später errechne und erstmal annehme, dass einer der Pulse = Takt ist. Ich habe das ganze jetzt mal auf einen langen Puls geändert. Ob das so wirklich besser ist weiss ich nicht.
Sorry, wir sind jetzt im try & error Mode

HomeAutoUser, 2017-11-07:
Hallo @sidey79 

>2017.11.07 20:31:38 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:10]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:32]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:24]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:40]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:103]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ: MC;P0=-597;P2=-378;P3=409;P4=623;D=35323232323232323232323232403232324032404032403232324032404040323232323232323232324032403232404032323232323232323232324040324040404040403;
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-597;LH=623;SL=-378;SH=409;D=DBAFFF5AAA8;C=334;L=41;R=243;
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:35]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:53]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:32]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:24]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:40]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:103]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ:  +seq[:36]
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ: MC;P0=-597;P2=-378;P3=409;P4=623;D=36;
2017.11.07 20:31:39 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-597;LH=623;SL=-378;SH=409;D=FFFBDADED5FFDBAFFF5AAA8;C=334;L=89;R=243;

>2017.11.07 20:34:22 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=395;P2=-10090;P3=264;P4=-390;P5=605;P6=-609;D=0123456341414141456141456145656141456141414141414141414561456141456561414141456141414145614141414565656565656;CP=1;SP=2;R=29;m=2;
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:14]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:41]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:56]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:61]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ:  +seq[:10]
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ: MC;P1=395;P4=-390;P5=605;P6=-609;D=10;
2017.11.07 20:34:22 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-609;LH=605;SL=-390;SH=395;D=FFFBBF76BBFF6EBEFBEAAA;C=333;L=87;R=29;
2017.11.07 20:34:33 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.07 20:34:33 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.07 20:34:33 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.07 20:34:33 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.07 20:34:33 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.07 20:34:33 4: Arduino_Nano_868Mhz/msg READ:  +seq[:41]
2017.11.07 20:34:33 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=397;P2=-399;P3=612;P4=-603;D=012121212121212121212121234121234121212121234121234123434121234123434343434123412123434121234343434121212341212341212123412123434341212;CP=1;R=29;

>2017.11.07 20:35:22 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:21]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:13]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:56]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:63]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ: MC;P3=-408;P4=599;P5=-623;P6=379;D=67636363636363636363636363456363634563454563456363634563454545636363636363636363634563456363454563636363636363636363634545634545454545456;
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-623;LH=599;SL=-408;SH=379;D=BAFFF5AAA8;C=334;L=37;R=245;
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:67]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:76]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:63]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:36]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:56]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ:  +seq[:60]
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ: MC;P3=-408;P4=599;P5=-623;P6=379;D=60;
2017.11.07 20:35:36 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-623;LH=599;SL=-408;SH=379;D=FFFBDADED5FFDBAFFF5AAA8;C=334;L=89;R=245;
2017.11.07 20:35:38 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READ:  +seq[:101]
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READ:  +seq[:12]
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READ:  +seq[:23]
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READ:  +seq[:34]
[... cut ...]
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READ:  +seq[:45]
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READ:  +seq[:56]
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READ:  +seq[:61]
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz/msg READredu: MU;P1=393;P4=-399;P5=610;P6=-613;D=171414141414141414141414145614141456145656561414561414141414145614141414141414141456145614145656141414561414145614141414561414565614561410;CP=1;R=7;
2017.11.07 20:35:40 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate

sidey79, 2018-02-18:
Ist das Problem noch vorhanden oder zwischenzeitlich gelöst?

elektron-bbs, 2018-02-18:
Dieses Problem besteht immer noch. Hier eine FS20- und eine FHT-Nachricht:

2018.02.18 17:38:08 4: sduino868IP/msg READ: Mc;LL=-610;LH=579;SL=-402;SH=366;D=FFF5D5F5DDFFF7BED58;C=326;L=73;R=245;
2018.02.18 17:38:08 4: sduino868IP/msg READ: MU;P0=387;P1=-404;P2=569;P3=-621;D=01012301012301010101010101010101010123010101230;CP=0;R=234;
2018.02.18 17:38:08 4: sduino868IP: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2018.02.18 17:38:08 4: sduino868IP: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2018.02.18 17:38:10 4: sduino868IP/msg READ: MU;P0=-160;P1=128;P2=-2464;P3=372;P4=-421;P5=578;P6=-605;P7=-9312;D=01234343434343434343434343456563434565656343434345656343456343456343434343434343434343434563434345634343434563456565634373434343434343434343434345656343456565634343434565634345634345634343434343434343434343456343434563434343456345656563437343434343434343;CP=3;R=243;O;
2018.02.18 17:38:10 4: sduino868IP: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2018.02.18 17:38:10 4: sduino868IP: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2018.02.18 17:38:10 4: sduino868IP: FS20 - remote control post demodulation 9C64000011 length 45
2018.02.18 17:38:10 4: sduino868IP: decoded matched MU Protocol id 74 dmsg 810b04f70101a0019C64000011 length 40 RSSI = -80.5
2018.02.18 17:38:10 4: sduino868IP: FS20 - remote control post demodulation 9C64000011 length 45
2018.02.18 17:38:10 4: sduino868IP: decoded matched MU Protocol id 74 dmsg 810b04f70101a0019C64000011 length 40 RSSI = -80.5
2018.02.18 17:38:10 4: sduino868IP Dispatch: 810b04f70101a0019C64000011, Dropped due to short time or equal msg
2018.02.18 17:38:10 4: sduino868IP/msg READ: MU;P0=-401;P1=399;P2=580;P3=-597;D=0101010101023101023232310101010232310102310102310101010101010;CP=1;R=217;
2018.02.18 17:38:10 4: sduino868IP: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2018.02.18 17:38:10 4: sduino868IP: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2018.02.18 17:38:11 4: sduino868IP/keepalive ok, retry = 0
2018.02.18 17:38:22 4: sduino868IP/msg READ: Mc;LL=-597;LH=585;SL=-395;SH=387;D=FFFB6EDEBAFFDAD6FFF6B6;C=327;L=87;R=33;
2018.02.18 17:38:22 4: sduino868IP/msg READ: MU;P1=583;P2=-401;P5=392;P6=-600;D=525252525252525252525252165216521652521652165252521616525216165252525252525252521652161652161652165252525252525252525252165216165216525;CP=5;R=33;
2018.02.18 17:38:22 4: sduino868IP: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2018.02.18 17:38:22 4: sduino868IP: FHT80 - roomthermostat post demodulation 52190000B600
2018.02.18 17:38:22 4: sduino868IP: decoded matched MU Protocol id 73 dmsg 810c04xx0909a00152190000B600 length 48 RSSI = -57.5
2018.02.18 17:38:22 4: sduino868IP: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2018.02.18 17:39:11 4: sduino868IP/keepalive ok, retry = 0

HomeAutoUser, 2018-02-18:
@elektron-bbs 
Generiere noch einmal deine Ausgabe mit aktivem DMC. Deine Zeiten bei MC sind definitiv verkehrt.
[Hier](https://github.com/RFD-FHEM/SIGNALDuino/issues/76#issuecomment-366534554) siehst du wie man DMC aktivieren kann in der Firmware. (signalDecoder.cpp)

elektron-bbs, 2018-02-18:
Das ist doch auch das Thema hier: "MU-Nachrichten werden z.T. als MC erkannt"
Deshalb habe ich eine FS20- und eine FHT-Nachricht gepostet, die fälschlicherweise auch als MC dekodiert werden.

sidey79, 2018-02-18:
@elektron-bbs 

Vielleicht stehe ich auf dem Schlauch... Ich habe diverse Signale gegen die aktuelle Lib getestet und konnte den Fehler nicht mehr nachstellen. Habe aber jetzt eine Signalfolge, mit der sich der Fehler nachstellen lässt. 
https://github.com/RFD-FHEM/SIGNALDuino/issues/76#issuecomment-366534554

sidey79, 2018-02-23:
Ich habe leider noch nichts finden können, wie ich dieses Fehlverhalten unterbinden, bin aber dran was zu suchen.

sidey79, 2018-02-24:
@elektron-bbs 

Ich denke ich habe einen Weg gefunden, diese Signale nicht mehr als MC zu erkennen.