RFD-FHEM/SIGNALDuino #75: MC - Decoder - OREGON kein Empfang mehr

HomeAutoUser, 2017-11-14:
Hallo @sidey79,
leider muss ich dir mitteilen, das der MC-Decoder einen Bug hat. In der[ aktuellen FW](https://github.com/RFD-FHEM/SIGNALDuino/tree/a03b1d6aefe0b81604023a3153759f3084e23c35) funktionieren **diese nicht mehr**. Ich musste [bis zur FW V 3.3.1-dev_171103](https://github.com/RFD-FHEM/SIGNALDuino/tree/2ddec7256416e618f9dda347df66a3f604bdfe8a)  downgraden das diese wieder funktionieren.
Anbei habe ich auch mal mich zurückgewagt bis zur [FW V 3.3.1-dev_170615](https://github.com/RFD-FHEM/SIGNALDuino/tree/1f56bbf9f300437567a78aecf457700f66a08aac) um diese gegen zu prüfen, weil dort auch die Hidekiprobleme wie [#hier schon mehrfach angeschrieben,](https://github.com/RFD-FHEM/SIGNALDuino/issues/68) nicht so gravieren waren.

Anbei eine Datei wo ich die Signale mal gegenüber gestellt habe.
Bitte diesem nachgehen, da sonst kein Empfang!

[OREGON_Probleme_MC.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/1472228/OREGON_Probleme_MC.txt)

sidey79, 2017-11-14:
@HomeAutoUser 

Ich kann deine Erkenntnisse nicht uneingeschränkt bestätigen. Habe hier auch ab und an mal einen OregonV1 Sensor den ich von den Nachbarn empfange. Gestern kam der mal.

Mit deinen Daten habe ich auch ein wenig getestet-

z.B.
```
V 3.3.1-dev_171107_1 SIGNALduino cc1101 - OREGON THR128 geht NICHT !!!!
https://github.com/RFD-FHEM/SIGNALDuino/tree/b48ed967c784ccbf28d21e225c6c52b2a6392023
MC;LL=-2761;LH=3110;SL=-1251;SH=1677;D=DFE1FFC1;C=1466;L=32;R=240; 
```
Geht bei mir ganz gut:

```
2017.11.14 22:07:08.231 4: THR128_04_1 decoded Oregon: T: 7.8 BAT: ok
2017.11.14 22:07:08.227 5: OREGON: checksum2 = 37 berechnet: 37
2017.11.14 22:07:08.226 5: OREGON: sensor_id=0a4d BitsMsg=80 Bits=80
2017.11.14 22:07:08.224 5: OREGON: decoding delay=199 hex=500A4D1004800700002500
2017.11.14 22:07:08.217 5: dummyDuino: dispatch 500A4D1004800700002500
2017.11.14 22:07:08.215 5: dummyDuino Dispatch: 500A4D1004800700002500, -82 dB, dispatch
2017.11.14 22:07:08.214 5: dummyDuino Dispatch: 500A4D1004800700002500, test ungleich: disabled
2017.11.14 22:07:08.212 4: dummyDuino: converted to hex: (500A4D1004800700002500)
2017.11.14 22:07:08.211 4: dummyDuino: OSV1 protocol id (18) translated to RFXSensor format
2017.11.14 22:07:08.208 4: dummyDuino: OSV1 input data: 00100000000111100000000000111110
2017.11.14 22:07:08.207 5: dummyDuino: extracted data 00100000000111100000000000111110 (bin)

2017.11.14 22:07:08.206 1: DEBUG>dummyDuino: extracted data 00100000000111100000000000111110 (bin)
2017.11.14 22:07:08.205 4: dummyDuino: Found manchester Protocol id 18 clock 1466 RSSI -82 -> OSV1
2017.11.14 22:07:08.204 1: DEBUG>clock and min length matched
2017.11.14 22:07:08.202 1: DEBUG>dummyDuino: processing manchester messag len:8

2017.11.14 22:07:08.200 1: DEBUG>dummyDuino: extracted RSSI 240 

2017.11.14 22:07:08.199 1: DEBUG>dummyDuino: extracted  number of 32 bits

2017.11.14 22:07:08.198 1: DEBUG>dummyDuino: extracted absolute clock 1466 

2017.11.14 22:07:08.197 1: DEBUG>dummyDuino: extracted  data DFE1FFC1

2017.11.14 22:07:08.195 1: DEBUG>dummyDuino: extracted  pattern SH 1677 

2017.11.14 22:07:08.194 1: DEBUG>dummyDuino: extracted  pattern SL -1251 

2017.11.14 22:07:08.193 1: DEBUG>dummyDuino: extracted  pattern LH 3110 

2017.11.14 22:07:08.191 1: DEBUG>dummyDuino: extracted  pattern LL -2761 

2017.11.14 22:07:08.190 1: DEBUG>dummyDuino: incomming message: (MC;LL=-2761;LH=3110;SL=-1251;SH=1677;D=DFE1FFC1;C=1466;L=32;R=240;)
```

Wichtig ist halt, dass Du nicht das FHEM Modul aus dem SVN verwendest, sonder aus dem dev-r33 Branch.

HomeAutoUser, 2017-11-14:
@sidey79,

>Wichtig ist halt, dass Du nicht das FHEM Modul aus dem SVN verwendest, sonder aus dem dev-r33 Branch.

[Ich verwende alles von hier.](https://github.com/RFD-FHEM/SIGNALDuino/tree/dev-r33_cc1101)

>Mit deinen Daten habe ich auch ein wenig getestet- ... Geht bei mir ganz gut:

Das ist richtig, DA HABE ICH MICH VERSSCHAUT, DANN probiere bitte **die u.g. Daten.**
Da erscheint keine Auswertung, ich habe diese nun soeben geschaut.

**Es fällt auch auf, sobald ich die FW wechsle, das die Erkennung keine Länge von L=32 erkennt.**
Dieser Sensor ist bei mir hier in der Wohnung, und wenn ich OHNE Änderung jeglicher Hardware oder Postitionen die FW downgrade funktioniert dieser. 

>V 3.3.1-dev_171114 SIGNALduino cc1101 - OREGON THR128 geht NICHT !!!!
https://github.com/RFD-FHEM/SIGNALDuino/tree/a03b1d6aefe0b81604023a3153759f3084e23c35
2017.11.14 19:30:47 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2735;LH=3155;SL=-1258;SH=1673;D=BE33C;C=1470;L=18;R=241;
2017.11.14 19:30:17 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2741;LH=3143;SL=-1243;SH=1669;D=BE33FE52;C=1465;L=31;R=3;
2017.11.14 19:29:48 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2725;LH=3147;SL=-1258;SH=1671;D=A7D3FF8C;C=1466;L=31;R=241;
2017.11.14 19:29:47 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2618;LH=3139;SL=-1260;SH=1676;D=A7D3FF8C;C=1448;L=31;R=240;
2017.11.14 19:28:18 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2623;LH=3189;SL=-1250;SH=1681;D=9B8FFAD0;C=1457;L=29;R=3;
2017.11.14 19:27:48 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2619;LH=3200;SL=-1247;SH=1684;D=A763FF34;C=1458;L=31;R=240;
2017.11.14 19:41:47 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2729;LH=3137;SL=-1257;SH=1676;D=BFB3FFD2;C=1466;L=31;R=241;
2017.11.14 19:42:17 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2738;LH=3152;SL=-1257;SH=1676;D=BFB3C;C=1470;L=18;R=239;
2017.11.14 19:43:17 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2730;LH=3131;SL=-1261;SH=1676;D=BFB3C;C=1466;L=18;R=241;
2017.11.14 19:45:17 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2713;LH=3149;SL=-1258;SH=1685;D=BEB3FED2;C=1467;L=31;R=242;
2017.11.14 19:45:17 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2714;LH=3148;SL=-1254;SH=1680;D=BEB3C;C=1465;L=18;R=241;
2017.11.14 19:46:47 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2729;LH=3146;SL=-1251;SH=1668;D=BEB3FED2;C=1465;L=31;R=240;
2017.11.14 19:47:17 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2713;LH=3148;SL=-1245;SH=1691;D=BF33C;C=1466;L=18;R=241;
2017.11.14 19:48:17 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2724;LH=3152;SL=-1246;SH=1677;D=BF33FF52;C=1466;L=31;R=240;
2017.11.14 19:48:47 4: Arduino_Nano_433Mhz/msg READ: MC;LL=-2664;LH=3155;SL=-1248;SH=1674;D=BF33FF52;C=1456;L=31;R=241;

sidey79, 2017-11-14:
Ich hatte wohl glück und genau den Teil erwischt der geklappt hat. :/
Kannst Du mir mal die Daten mit deaktivierem MC Decoder als MU Nachrichten senden?

Ich glaube dir ja dass es an der FW liegt. Ich finde nur den oder die Fehler im Zusammenhang mit dem MC nicht. :(

HomeAutoUser, 2017-11-14:
Hallo @sidey79,

ich habe nun mal einige Zeit dies als MU mitloggen lassen. Kannst du die parallel dazu mal die Patternausgabe ansehen. Diese kommt auch ab und zu in der Form.
>2017.11.14 23:10:47 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-27224;P1=1673;P2=-1260;P3=-4304;P4=5712;P5=-6752;P6=3145;P7=-2718;D=012121212121212121212121345126712671212121267621712121212121212121212621712126;CP=1;R=245;
2017.11.14 23:10:47 4: Arduino_Nano_433Mhz/keepalive ok, retry = 0
2017.11.14 23:10:48 4: Arduino_Nano_433Mhz/msg READ: after compress_pattern ->
2017.11.14 23:10:48 4: Arduino_Nano_433Mhz/msg READ:  mstart=0 mend=0 msglen=70 bytecnt=34 valcnt=70 mTrunc=0 state=3 wrong Data in Buffer
2017.11.14 23:10:48 4: Arduino_Nano_433Mhz/msg READ: 
2017.11.14 23:10:48 4: Arduino_Nano_433Mhz/msg READ: Sync:  Clock: NULL, Tol: 632, PattLen: 8 (0), Pulse: -32001, 3136, mStart: 0, MCD: 0, mtrunc: 0
2017.11.14 23:10:49 4: Arduino_Nano_433Mhz/msg READ: Signal: 0101010101010101345106710671010101067601710101010101010101010601710106.  [70]
2017.11.14 23:10:49 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 28*[1099] P1: 28*[1676] P2: 0*[] P3: 1*[-4300] P4: 1*[5704] P5: 1*[-6760] P6: 6*[3136] P7: 5*[-2720]
2017.11.14 23:10:49 4: Arduino_Nano_433Mhz/msg READ: MU;P0=1099;P1=1676;P3=-4300;P4=5704;P5=-6760;P6=3136;P7=-2720;D=0101010101010101345106710671010101067601710101010101010101010601710106;CP=0;R=3;

>2017.11.14 23:13:32 4: Arduino_Nano_433Mhz/msg READ: MU;P0=196;P1=-1023;P2=914;P3=436;P4=-1531;P5=1401;P6=-542;P7=-17944;D=012121212121345656563456565634343456345634343434563456272121212121212121212121212121212121213456565634565656343434563456343434345634562;CP=2;R=213;
2017.11.14 23:13:35 4: Arduino_Nano_433Mhz/msg READ: after compress_pattern ->
2017.11.14 23:13:35 4: Arduino_Nano_433Mhz/msg READ:  mstart=0 mend=0 msglen=254 bytecnt=126 valcnt=254 mTrunc=0 state=3 wrong Data in Buffer
2017.11.14 23:13:35 4: Arduino_Nano_433Mhz/msg READ: 
2017.11.14 23:13:35 4: Arduino_Nano_433Mhz/msg READ: Sync:  Clock: NULL, Tol: 208, PattLen: 8 (1), Pulse: 908, -1043, mStart: 0, MCD: 0, mtrunc: 0
2017.11.14 23:13:35 4: Arduino_Nano_433Mhz/msg READ: Signal: 02323232323232323232323232323232345254345254345254345254345232543234523254323232345202540452540452540204520254045254020452020202025404520202540204520202020202020202025404520254020204520202540202020204525404525404525404520254045254045202670202020202020212.  [254]
2017.11.14 23:13:35 4: Arduino_Nano_433Mhz/msg READ: Pattern:  P0: 55*[-279] P1: 0*[] P2: 83*[-1043] P3: 28*[938] P4: 43*[-527] P5: 43*[428] P6: 1*[112] P7: 1*[-2256]
2017.11.14 23:13:35 4: Arduino_Nano_433Mhz/msg READ: MU;P0=423;P1=-1050;P2=-526;P3=929;D=01023201023201310231320131023131313201310232010;CP=0;R=219;

**MU Ausgaben THR128** (ich hoffe die richtigen Zeiten erkannt zu haben)
>2017.11.14 23:10:47 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-27224;P1=1673;P2=-1260;P3=-4304;P4=5712;P5=-6752;P6=3145;P7=-2718;D=012121212121212121212121345126712671212121267621712121212121212121212621712126;CP=1;R=245;
2017.11.14 23:10:49 4: Arduino_Nano_433Mhz/msg READ: MU;P0=1099;P1=1676;P3=-4300;P4=5704;P5=-6760;P6=3136;P7=-2720;D=0101010101010101345106710671010101067601710101010101010101010601710106;CP=0;R=3;
2017.11.14 23:11:17 4: Arduino_Nano_433Mhz/msg READ: MU;P0=1012;P1=-1253;P2=1678;P3=-4276;P4=5724;P5=-6752;P6=3154;P7=-2711;D=012121212121212123452167216721612127216127212121212121212161276721216;CP=2;R=242;
2017.11.14 23:11:19 4: Arduino_Nano_433Mhz/msg READ: MU;P0=968;P1=-1255;P2=1675;P3=-4280;P4=5736;P5=-6736;P6=3134;P7=-2723;D=012121212121212123452167216721612127216127212121212121212161276721216;CP=2;R=242;
2017.11.14 23:11:47 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-2721;P1=1680;P2=-1261;P3=-4264;P4=5740;P5=-6736;P6=3140;D=0121212121212121213451260126012621210126210121212121212121262106012126;CP=1;R=3;
2017.11.14 23:11:48 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-2714;P1=1687;P2=-1250;P3=-4280;P4=5728;P5=-6724;P6=3143;D=0121212121212121213451260126012621210126210121212121212121262106012126;CP=1;R=245;
2017.11.14 23:12:17 4: Arduino_Nano_433Mhz/msg READ: MU;P0=1208;P1=-1243;P2=1684;P3=-4280;P4=5716;P5=-6748;P6=3153;P7=-2728;D=0121212121212123452167216721612127216127212121212121212161276721216;CP=2;R=244;
2017.11.14 23:12:19 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-2716;P1=1682;P2=-1252;P3=-4300;P4=5708;P5=-6764;P6=3143;D=01212121212121213451260126012621210126210121212121212121262106012126;CP=1;R=245;
2017.11.14 23:12:47 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-2723;P1=1256;P2=-1255;P3=1675;P4=-4292;P5=5700;P6=-6756;P7=3153;D=12323232323234563270327032723230327230323232323232323272307032327;CP=3;R=198;

sidey79, 2017-11-15:
Das sieht brauchbar aus. Schaue ich mir an :)

HomeAutoUser, 2017-11-15:
@sidey79 
Da mit deiner Modifizierung der OREGON wieder empfangbar ist und auch die Hideki Sesnoren ein positives Verhalten aufweisen fürs erste, so würde ich den Faden schließen.

sidey79, 2017-11-15:
Ist dass jetzt so gut wie in etwa im Juli mit der FW?

HomeAutoUser, 2017-11-15:
Was ich auf den ersten Blick erkennen kann sieht es gut aus. Für eine abschließende Aussage müsste ich mal speziell ein Logfile anlegen mit den neu eingebauten MSG_Counter Reading vom Hideki. So sehen wir ob alle Nachrichtenteile eintreffen. So siehst du auch selbst das Verhalten mal vor Augen. Es wäre schön, wenn @Ralf9 das vielleicht ebenso machen könnte, um diese Erkenntnis von einem LowBudget Empfänger + CC1101 von meiner Seite zu haben.

So sehen wir im Logfile Bsp. im Optimalfall: (Weil max. 3 Wiederholungen im Signal sind)
Zeit 1: RAW_MSG (Teil1)
Zeit 1: Counter1
Zeit 1: RAW_MSG (Teil2)
Zeit 1: Counter2
Zeit 1: RAW_MSG (Teil3)
Zeit 1: Counter3
Zeit 2: RAW_MSG (Teil1)
Zeit 2: Counter1
Zeit 2: RAW_MSG (Teil2)
Zeit 2: Counter2
Zeit 2: RAW_MSG (Teil3)
Zeit 2: Counter3
.
.
.

sidey79, 2017-11-15:
Prima. Ich habe mich nun (obwohl schon geschlossen) noch um diese Preamble Sequenz gekümmert.
Damit wird der Sensor auch richtig dekodiert, wenn die Preamble Sequenz mit empfangen wurde :)

Nebenbei habe ich noch ein paar der Intinsiven Abfragen gegen den Puffer abgefangen.

HomeAutoUser, 2017-12-04:
Hallo @Ralf9 , @sidey79 

Bei mir hatte seit einigen Tagen bzw. Commit´s ein Sensor ein "unauffäliges Verhalten".
Er bekam ständig Spitzen obwohl was anderes auf dem Display stand.

>2017-12-04_20:33:24 THR128_1 T: 8.1 BAT: ok
2017-12-04_20:33:24 THR128_1 RSSI: -94
2017-12-04_20:33:24 THR128_1 DMSG: 500A4D1004100800001F00
2017-12-04_20:33:54 THR128_1 T: 8.1 BAT: ok
2017-12-04_20:33:54 THR128_1 RSSI: -50.5
2017-12-04_20:33:54 THR128_1 DMSG: 500A4D1004100800001F00
2017-12-04_20:34:24 THR128_1 T: 8.2 BAT: ok
2017-12-04_20:34:24 THR128_1 DMSG: 500A4D1004200800002000
2017-12-04_20:34:24 THR128_1 RSSI: -79.5
2017-12-04_20:34:54 THR128_1 T: 4.1 BAT: ok
2017-12-04_20:34:54 THR128_1 RSSI: -79
2017-12-04_20:34:54 THR128_1 DMSG: 500A4D1002100400001900
2017-12-04_20:37:24 THR128_1 T: 4.1 BAT: ok
2017-12-04_20:37:24 THR128_1 RSSI: -79
2017-12-04_20:37:24 THR128_1 DMSG: 500A4D1002100400001900
2017-12-04_20:40:24 THR128_1 T: 7.9 BAT: ok
2017-12-04_20:40:24 THR128_1 DMSG: 500A4D1004900700002600
2017-12-04_20:40:24 THR128_1 RSSI: -79.5
2017-12-04_20:40:54 THR128_1 T: 7.9 BAT: ok
2017-12-04_20:40:54 THR128_1 DMSG: 500A4D1004900700002600
2017-12-04_20:40:54 THR128_1 RSSI: -79

>2017-12-04_21:19:24 THR128_1 T: 8.1 BAT: ok
2017-12-04_21:19:24 THR128_1 RSSI: -95
2017-12-04_21:19:24 THR128_1 DMSG: 500A4D1004100800001F00
2017-12-04_21:20:24 THR128_1 T: 4 BAT: ok
2017-12-04_21:20:24 THR128_1 RSSI: -80
2017-12-04_21:20:24 THR128_1 DMSG: 500A4D100A000400002000
2017-12-04_21:20:54 THR128_1 T: 8 BAT: ok
2017-12-04_21:20:54 THR128_1 RSSI: -80
2017-12-04_21:20:54 THR128_1 DMSG: 500A4D1004000800001E00

Ein doubleCheckMsg bringt in dem Falle nichts, was ich schon testete. Könnt Ihr euch das erklären ?
@Ralf9, bei deiner FW ist dies auch der Fall.

Wenn das bedingt wegen dem "schlechten Empfang" ist (obwohl der Sender ca. aller 60Sek. Werte liefert) , so würde ich vorschlagen bzw. auch favourisieren in das [Modul OREGON](https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/41_OREGON.pm) einen Sani Check einzuarbeiten. Diese "Spitzen" würden dann somit geglättet werden.

sidey79, 2017-12-04:
Seltsam, der CRC Check scheint ja das Problem nicht zu finden... Wie gut der crc beim osv1 nun ist, habe ich aber auch nicht verifiziert.. 

Ist ein bisschen tricky im oregon modul, den Plaulibilitätscheck bräuchte man ja nur auf diesen sensor nehme ich an.

HomeAutoUser, 2017-12-04:
Ich habe nun das ganze mal hin und her getestet. Wenn ich zurück zum [Commit](https://github.com/RFD-FHEM/SIGNALDuino/tree/6414e4fbd60901ee51e61e3c0fefb7ed2be7bae6) - dev_r33_cc1101 gehe scheint dies zu funktionieren.

Sobald ich den [1. Commit](https://github.com/RFD-FHEM/SIGNALDuino/tree/e0d0897d56d6a2c660a2c09521d122c4fd4657c4) dev_r33_cc110_mc_firtbitfix einspiele, so fangen die Probleme an. Im Log äußert sich auch dies so,
>2017.12.04 22:19:41 3: Arduino_Nano_433Mhz/init: enable receiver (XE)
2017.12.04 22:19:54 3: OREGON: Unknown device THR128_4, please define it
2017.12.04 22:20:24 3: OREGON: Unknown device THR128_4, please define it

und der "herkömmliche" Sensor wird nicht mehr empfangen. Gehe ich eine - 2 Commit´s höher so empfange ich den Sensor aber mit den Spitzen wie oben schon erwähnt.

>Ist ein bisschen tricky im oregon modul, den Plaulibilitätscheck bräuchte man ja nur auf diesen sensor nehme ich an.

Ja ist eine Variante. Wir sollten nur überlegen, wenn wir so einen Saniy Check einmal in die "Hand nehmen", ob wir diesen dann nicht gleich "komplett" umsetzen wollen im Modul. So würden wir es komplett vollenden und nicht nur "Stückchen" anfassen.

sidey79, 2017-12-04:
Es ist echt verhext..

Will ich was richte, gibt es immer neu Probleme :(

HomeAutoUser, 2017-12-04:
*Kopf hoch*
Es wird einfach daran liegen das es sehr viele Baustellen sind auf einmal. Neben den Verbesserungen und Fixen sollen wir vielleicht versuchen erstmal an Dingen dran zu bleiben wo Fehler sind und diese versuchen zu finden. Es ist kein Rückschlag der Bug. Er wurde entdeckt und deshalb befinden wir uns auch in DEV! Desto mehr Änderungen auf einmal, desto mehr "verwirbeln" wir und verschleiern womöglich Fehler.

HomeAutoUser, 2017-12-04:
@sidey79 ich habe vielleicht einen Ansatz gefunden woran es liegt. Schaue dir mal folgendes an.

funktionierende Nachricht!
>2017.12.04 23:25:54 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2761;LH=3111;SL=-1283;SH=1642;D=2C8100AD;C=1466;L=32;R=242;
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1466 RSSI -81 -> OSV1
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: OSV1 input data: 00101100100000010000000010101101
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: converted to hex: (500A4D1004100800001F00)
2017.12.04 23:25:54 3: OREGON: decoding delay=60 hex=500A4D1004100800001F00
2017.12.04 23:25:54 3: OREGON: sensor_id=0a4d BitsMsg=80 Bits=80
2017.12.04 23:25:54 3: OREGON: checksum2 = 31 berechnet: 31
2017.12.04 23:25:54 3: THR128_1 decoded Oregon: T: 8.1 BAT: ok
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2657;LH=3117;SL=-1290;SH=1732;D=2C8100AD;C=1465;L=32;R=241;
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1465 RSSI -81.5 -> OSV1
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: OSV1 input data: 00101100100000010000000010101101
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz: converted to hex: (500A4D1004100800001F00)
2017.12.04 23:25:54 4: Arduino_Nano_433Mhz Dispatch: 500A4D1004100800001F00, Dropped due to short time or equal msg

FALSCHE NACHRICHT!
>2017.12.04 23:23:54 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2740;LH=3126;SL=-1276;SH=1661;D=2C41006D;C=1467;L=32;R=243;
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1467 RSSI -80.5 -> OSV1
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: OSV1 input data: 00101100010000010000000001101101
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: converted to hex: (500A4D1004200800002000)
2017.12.04 23:23:54 3: OREGON: decoding delay=30 hex=500A4D1004200800002000
2017.12.04 23:23:54 3: OREGON: sensor_id=0a4d BitsMsg=80 Bits=80
2017.12.04 23:23:54 3: OREGON: checksum2 = 32 berechnet: 32
2017.12.04 23:23:54 3: THR128_1 decoded Oregon: T: 8.2 BAT: ok
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz/msg READredu: MU;P0=348;P2=-25648;P3=506;P4=-2109;P5=-4144;P6=-9032;P7=-812;D=23435353435343535343534343634363535343437;CP=3;R=8;e;
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2332;LH=3122;SL=-1277;SH=1746;D=588200DA;C=1412;L=31;R=243;
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1412 RSSI -80.5 -> OSV1
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: OSV1 input data: 01011000100000100000000011011010
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.04 23:23:54 4: Arduino_Nano_433Mhz: converted to hex: (500A4D100A100400002100)
2017.12.04 23:23:54 3: OREGON: decoding delay=0 hex=500A4D100A100400002100
2017.12.04 23:23:54 3: OREGON: sensor_id=0a4d BitsMsg=80 Bits=80
2017.12.04 23:23:54 3: OREGON: checksum2 = 33 berechnet: 33
2017.12.04 23:23:54 3: THR128_1 decoded Oregon: T: 4.1 BAT: ok

Wenn ich nun beide input data übereinander stelle
00_101100010000010000000001101101 T: 8.2 BAT: ok
0__101100010000010000000001101101_0 T: 4.1 BAT: ok
fällt dir was auf? Es liegt an der Erkennung bzw MC-Trennung.
Der Sensor wird auch mehrfach senden. Ich vermute 2x und so wird durch eine falsche Trennung der Wert Verschoben. --->> MC - Trennung Baustelle.

sidey79, 2017-12-05:
Sehr guter Hinweis.

Das könnte daran liegen, dass der Sync Puls nicht im Puffer war oder anderweitig verworfen wurde.

Ich werde mal testen was passiert, wenn ich das Signal mehrfach in den Decoder schiebe....

sidey79, 2017-12-05:
@HomeAutoUser 

Ich habe mir das mal näher angesehen.
Mit commit e0d0897d56d6a2c660a2c09521d122c4fd4657c4 hat sich tatsächlich was geändert. Das Stimmt. Die Daten sind invertiert

Und ja, das Problem ist, es fehlt uns ein Bit in der Übertragung. Der CRC ist dafür zu simpel.

Wir könnten Nachrichten mit 31 Bits ignorieren, da diese nicht vollständig sind. Das Löst das Problem in FHEM, aber nicht vom Empfang.

Ich habe Folgende Daten 2x in den Decoder geschickt:

`MU;P0=-27224;P1=1673;P2=-1260;P3=-4304;P4=5712;P5=-6752;P6=3145;P7=-2718;D=012121212121212121212121345126712671212121267621712121212121212121212621712126;CP=1;R=245;`

bis commit e0d0897d56d6a2c660a2c09521d122c4fd4657c4 kam dabei heraus:
DBE9FFCE
11011011111010011111111111001110

Jetzt ist es
24160031
00100100000101100000000000110001

Das Signal bekomme ich auch 2x extrahiert. Ich weiss jetzt nicht exakt, wie der Oregon Sensor seine Wiederholung trennt.

HomeAutoUser, 2017-12-06:
@sidey79 
ich habe mir mal mit viel Mühe die Arbeit gemacht und versucht mitzuschneiden was der SIGNALduino bei ausgeschaltenem MC erkennt.

```
2017.12.06 12:38:54 4: radino_Stick_433Mhz/msg READredu: MU;P0=-2745;P1=-13888;P2=5444;P3=-1288;P4=1642;P5=-4304;P6=-6768;P7=3112;D=2343434343434343434343452643704343434343437043434343437043434343434370704343434;CP=4;R=248;e;
2017.12.06 12:38:54 4: radino_Stick_433Mhz/msg READredu: MU;P0=-27304;P1=-1290;P2=1632;P3=-4272;P4=5680;P5=-6776;P6=3101;P7=-2760;D=12121212121212121212123452167212121212121672121212121672121212121216767212121200;CP=2;R=221;
```

```
2017.12.06 13:19:24 4: radino_Stick_433Mhz/msg READredu: MU;P0=-16360;P1=-1286;P2=1641;P3=-4320;P4=5680;P5=-6752;P6=3104;P7=-2747;D=12121212121212121212123452167212121216127212121212121672121212121672167212121200;CP=2;R=252;
2017.12.06 13:19:25 4: radino_Stick_433Mhz/msg READredu: MU;P0=32001;P1=-1301;P2=1634;P3=-4304;P4=5688;P5=-6760;P6=3110;P7=-2747;D=01212121212121212121212345216721212121612721212121212167212121212167216721212120;CP=2;R=219;e;
```

```
2017.12.06 13:14:54 4: radino_Stick_433Mhz/msg READredu: MU;P0=4192;P1=-1302;P2=1620;P3=-4280;P4=5712;P5=-6760;P6=3106;P7=-2753;
D=01212121212121212121212345216721212121612721212121212167212121212167216721212120;CP=2;R=248;e;
2017.12.06 13:14:54 4: radino_Stick_433Mhz/msg READredu: MU;P0=3213;P1=-1307;P2=1627;P3=-4296;P4=5696;P5=-6744;P7=-2765;D=01212121212121212121212345210721212121012721212121212107212121212107210721212120;CP=2;R=247;e;
```

Vielleicht kannst du anhand dieser Ausgaben neue Erkenntnisse für dich herbei ableiten.

sidey79, 2017-12-06:
Ist das mit der Firmware aus diesem Repo?

Die Ausgabe e am Ende kenne ich nicht. Wichtig wäre halt, dass ich alle Daten aus dem Message Puffer habe. Wenn es sich hier um den Fork von Ralf handelt, dann wird bei MU nicht mehr der komplette Signalpuffer ausgegeben. Zumindest so mein Verständnis.

Ich werde die MU Nachrichten in den Decoder jagen, kann aber gut sein, dass der Fehler so nicht zu finden ist. Vor allem befinden sich Signalfolgen die nicht valide sind.

Bitte nicht falsch verstehen, es ist nur verdammt schwer einen Fehler zu finden, wenn man nicht weiss wie es zu diesen MU Daten kommt.

Ralf9, 2017-12-06:
Das "e" bedeutet, daß ein Nachrichtenende erkannt wurde.
Bei einem "w" am Ende wurde eine Pause erkannt (MUcut) und es wird nur bis zur erkannten Pause ausgegeben.

Ich habe mir mal die Oregon Protokollbeschreibung angesehen:
```
Version 1.0 sensors have a simpler format as shown in figure 2 below.
1. The preamble contains twelve “1” bits.
2. The sync section consists of a long off period (4.2msec), a long RF pulse
(5.7msec) and another long off period (around 5msec).
3. The first data sample point (clock edge) is not always marked by an RF
transition and must be measured from the end of the long sync pulse.
```
Dies ist der Sync
P3=-4320;P4=5680;P5=-6752;D=345
je nachdem das erste Datenbit 0 oder 1 ist, ist P5 5000-5400 oder 6480-6800

Diese Abhängigkeit von P5 dürfte es sehr schwierig machen, das erste Datenbit sauber zu erkennen.
Bei dem Somfy Protokoll besteht ein ähnliches Problem.

sidey79, 2017-12-06:
Der Sync wird berücksichtigt, das hatte ich ja erst kürzlich behoben.

sidey79, 2017-12-06:
Ich habe mir erst mal was basteln müssen, damit ich etwas komfortabler Daten verifizieren kann :)

Wenn ich dies Daten in den Decoder einlese:
` MU;P0=-2745;P1=-13888;P2=5444;P3=-1288;P4=1642;P5=-4304;P6=-6768;P7=3112;D=2343434343434343434343452643704343434343437043434343437043434343434370704343434;
`

Habe ich folgendes im Puffer: 
```
Sync:  Clock: 1642, Tol: 328, PattLen: 7 (7), Pulse: 1642, -1288, mStart: 26, MCD: 0, mtrunc: 0
Signal: 0121212121212121212121230421562121212121215621212121215621212121212156562121212.  [79]
Pattern:  P0: 2*[5444] P1: 32*[-1288] P2: 33*[1642] P3: 1*[-4304] P4: 1*[-6768] P5: 5*[3112] P6: 5*[-2745]
```

Und der MC Decoder gibt folgende Debug  Meldung aus:
`DMc;P1=-1288;P2=1642;P3=-4304;D=
`
Das finde ich seltsam, habe ich aber nicht untersucht, wieso hier die Ausgabe der Daten nicht klappt. Scheint ein Fehler zu sein.

Und Folgende Daten erhalte für die Manchester Nachricht
`Mc;LL=-2745;LH=3112;SL=-1288;SH=1642;D=20204050;C=1464;L=32;R=0;

Im Puffer sind dann immer noch Daten, die sich nicht mehr als MC Signal erkennen lassen.
```
Sync:  Clock: 1642, Tol: 328, PattLen: 7 (7), Pulse: 1642, -1288, mStart: 26, MCD: 0, mtrunc: 0
Signal: 0121212121212121212121230421562121212121215621212121215621212121212156562121212.  [79]
Pattern:  P0: 2*[5444] P1: 32*[-1288] P2: 33*[1642] P3: 1*[-4304] P4: 1*[-6768] P5: 5*[3112] P6: 5*[-2745]
```

Ich mach für heute erst mal Schluss

HomeAutoUser, 2017-12-07:
Hallo @sidey79 
schau mal hier, [vielleicht hilft dir dies etwas weiter (ab Zeile 333) ](https://github.com/merbanan/rtl_433/blob/master/src/pulse_demod.c) bzw. du kannst daraus etwas nützliches abschauen. Hier wurde auf das Problem mit ACHTUNG hingewiesen. Er scheint dies auch separat zu behandeln.

>/*
 * Oregon Scientific V1 Protocol
 * Starts with a clean preamble of 12 pulses with
 * consistent timing followed by an out of time Sync pulse.
 * Data then follows with manchester encoding, but
 * care must be taken with the gap after the sync pulse since it
 * is outside of the normal clocking.  Because of this a data stream
 * beginning with a 0 will have data in this gap.
 * This code looks at pulse and gap width and clocks bits
 * in from this.  Since this is manchester encoded every other
 * bit is discarded.
 */

Hier kann mal auch [einiges erlesen über die Pulse detection functions](https://github.com/merbanan/rtl_433/blob/master/src/pulse_detect.c) bzw. [hier über die Pulse demodulation functions](https://github.com/merbanan/rtl_433/blob/master/include/pulse_demod.h). Vielleicht kann dies sehr nützlich sein.

sidey79, 2017-12-07:
Sieht vielversprechend aus. Ich schau mir das Mal an

sidey79, 2017-12-07:
@HomeAutoUser 
Ich habe mir das angesehen. Ich weiss nicht nicht so ganz was für Daten dort verarbeitet werden

Da wird bit 1 und 0 fest angegebem
https://github.com/merbanan/rtl_433/blob/4a2f763a9349164b7ec20bf90fb0b9d899e1f483/src/pulse_demod.c#L252

https://github.com/merbanan/rtl_433/blob/4a2f763a9349164b7ec20bf90fb0b9d899e1f483/src/pulse_demod.c#L276

Ich verstehe aber nicht, wie erkannt wird ob es eine high oder ein low pegel ist.

Vielleicht ist das 
pulses->pulse
und 
pulses->gap
??

HomeAutoUser, 2017-12-08:
@sidey79 

Wenn ich den Code richtig deute, _so entnehme ich dem, das bei Oregon Sensoren des etwas spezielles ist,_ als andere Sensoren mit MC. Er schreibt auch [hier](https://github.com/merbanan/rtl_433/blob/4a2f763a9349164b7ec20bf90fb0b9d899e1f483/src/pulse_demod.c#L244) das dies es eine Eigenheit von denen sein wird, sozusagen "Sonderlinge im MC Verfahren" ;-)

Darauf aufbauend [wird bit 1](https://github.com/merbanan/rtl_433/blob/4a2f763a9349164b7ec20bf90fb0b9d899e1f483/src/pulse_demod.c#L252) gesetzt. 
>Das letzte Bit wurde mehr als [short_limit](https://github.com/merbanan/rtl_433/blob/master/src/devices/oregon_scientific_v1.c#L85) * 1.5 Mal aufgenommen

... // also muss dieser Pulsstart eine Datenflanke sein (fallende Datenflanke bedeutet Bit = 1)

sidey79, 2017-12-08:
Hallo @homeautouser

Ich habe die grundsätzliche Funktionsweise noch nicht komplett verstanden.

Ich nehme an, dass im Array Pulse die High Pegel Zeiten erfasst sind und im Array Gap die Low Pegel.

Angefangen wird immer mit einem High Pegel, das erscheint sinnvoll :)

Erst Mal möchte ich den Manchester Decoder erkennen. Dort werden die Zeiten der Pulse ausgewertet. Sieht so aus, als ob einfach die verstrichene Zeit in einer Variable durch Addition gespeichert wird.

Ob das Verfahren nun besser als das von mir entwickelte ist weiss ich nicht. 

Was ich halt zunächst nicht gesehen hatte ist, dass in einem Seifendurchlauf zwei Werte verarbeitet werden, da zwei Arrays verwendet werden.

HomeAutoUser, 2017-12-09:
@sidey79 
ich werde die [von dir geänderte FW](https://github.com/RFD-FHEM/SIGNALDuino/tree/b258f548ec37d7dc73c92344490b538d11a5069f) mal testen. Sollte der Fehler wieder auftreten so werde ich mal in den Apfel beißen und solange mitloggen mit vebose 5 bis dies auftritt.

Mit der derzeitigen FW wo der Fehler auftritt habe ich die Nacht völlig falsche werte erhalten.
>2017-12-09_12:44:55 THR128_4 T: -159.4 BAT: low
2017-12-09_12:44:55 THR128_4 DMSG: 500A4D400BE4F808005100
2017-12-09_12:44:55 THR128_4 RSSI: -87.5
2017-12-09_12:44:55 THR128_4 RAWMSG: Mc;LL=-2628;LH=3212;SL=-1257;SH=1664;D=D371FF5A;C=1460;L=32;R=229;
2017-12-09_07:27:55 THR128_4 temperature: -159.2
2017-12-09_07:27:55 THR128_4 battery: low
2017-12-09_07:27:55 THR128_4 T: -159.2 BAT: low
2017-12-09_07:27:55 THR128_4 RAWMSG: Mc;LL=-2693;LH=3163;SL=-1229;SH=1704;D=D331FF1A;C=1464;L=32;R=253;
2017-12-09_07:27:55 THR128_4 RSSI: -75.5
2017-12-09_07:27:55 THR128_4 DMSG: 500A4D400BC4F808004F00

Weitere Erkenntnisse werde ich Euch berichten.

Ralf9, 2017-12-09:
> die wenigen Minuten wo ich die FW von Dir seoben laufen lassen habe, empfange ich nun den Sensor garnicht mehr. Dafür deutet er öfter nun wie o.genannt schon

Dieses Signal ist invertiert.
```
2017-12-09_13:29:55 THR128_4 RAWMSG: Mc;LL=-2696;LH=3164;SL=-1244;SH=1683;D=DB31FF16;C=1464;L=32;R=231;
2017.12.09 13:42:42.633 4 : THR128_1 decoded Oregon: T: 7.3 BAT: ok
```

Als vorläufigen Workaround kannst Du Dir mit einem zusätzlichem Protokoll 18.1 helfen:

```
	"18.1"    => 			## Oregon Scientific v1
		{
            name			=> 'OSV1',	
			id          	=> '18',
			clockrange     	=> [1400,1500],			# min , max
			format 			=> 'manchester',	    # tristate can't be migrated from bin into hex!
			preamble		=> '',					
			clientmodule    => 'OREGON',
			modulematch     => '^[0-9A-F].*',
			length_min      => '31',
			length_max      => '32',
			#polarity        => 'invert',		    # invert bits
			method          => \&SIGNALduino_OSV1   # Call to process this message
		},
```

HomeAutoUser, 2017-12-09:
Ich habe mir den Workaround vorrübergehend eingerichtet um auch Erkenntnisse zu sammeln.
So wie es aussieht, Empfange ich nun den Sensor auf 2 Varianten. Einen Teil der MSG als "normal" den anderen als "invertiert".

Auszug Readingsgroup der Empfänge (Gegenüberstellung Protokoll 18 zu 18.1:
>Küche_Kühlschrank | 2017-12-09 13:54:55 | -77 dB | 8
Küche_Kühlschrank_invertiert | 2017-12-09 13:54:55 | -77 dB | 8

HomeAutoUser, 2017-12-09:
Korrekter weise hier die Erkenntnisse aus dem versehendlichen falschen Faden.

@sidey79, die wenigen Minuten wo ich die FW von Dir seoben laufen lassen habe, empfange ich nun den Sensor garnicht mehr. Dafür deutet er nur noch wie o.genannt schon den Sensor wie folgt.

>2017-12-09_13:29:55 THR128_4 temperature: -159.2
2017-12-09_13:29:55 THR128_4 battery: low
2017-12-09_13:29:55 THR128_4 T: -159.2 BAT: low
2017-12-09_13:29:55 THR128_4 DMSG: 500A4D400BC4F808004F00
2017-12-09_13:29:55 THR128_4 RAWMSG: Mc;LL=-2696;LH=3164;SL=-1244;SH=1683;D=DB31FF16;C=1464;L=32;R=231;
2017-12-09_13:29:55 THR128_4 RSSI: -86.5
2017-12-09_13:30:55 THR128_4 temperature: -159.1
2017-12-09_13:30:55 THR128_4 battery: low
2017-12-09_13:30:55 THR128_4 T: -159.1 BAT: low
2017-12-09_13:30:55 THR128_4 RSSI: -78.5
2017-12-09_13:30:55 THR128_4 DMSG: 500A4D400BB4F808004E00
2017-12-09_13:30:55 THR128_4 RAWMSG: Mc;LL=-2729;LH=3141;SL=-1254;SH=1683;D=D3D1FFEA;C=1467;L=32;R=247;
2017-12-09_13:31:55 THR128_4 temperature: -159.1
2017-12-09_13:31:55 THR128_4 battery: low
2017-12-09_13:31:55 THR128_4 T: -159.1 BAT: low
2017-12-09_13:31:55 THR128_4 RSSI: -81
2017-12-09_13:31:55 THR128_4 DMSG: 500A4D400BB4F808004E00
2017-12-09_13:31:55 THR128_4 RAWMSG: Mc;LL=-2699;LH=3158;SL=-1245;SH=1682;D=D3D1FFEA;C=1463;L=32;R=242;
2017-12-09_13:32:25 THR128_4 temperature: -159.1
2017-12-09_13:32:25 THR128_4 battery: low
2017-12-09_13:32:25 THR128_4 T: -159.1 BAT: low
2017-12-09_13:32:25 THR128_4 RAWMSG: Mc;LL=-2707;LH=3154;SL=-1262;SH=1673;D=D3D1FFEA;C=1465;L=32;R=245;
2017-12-09_13:32:25 THR128_4 DMSG: 500A4D400BB4F808004E00
2017-12-09_13:32:25 THR128_4 RSSI: -79.5
2017-12-09_13:32:55 THR128_4 temperature: -159
2017-12-09_13:32:55 THR128_4 battery: low
2017-12-09_13:32:55 THR128_4 T: -159 BAT: low
2017-12-09_13:32:55 THR128_4 DMSG: 500A4D400BA4F808004D00
2017-12-09_13:32:55 THR128_4 RAWMSG: Mc;LL=-2726;LH=3144;SL=-1264;SH=1671;D=D351FF6A;C=1467;L=32;R=247;
2017-12-09_13:32:55 THR128_4 RSSI: -78.5
2017-12-09_13:33:25 THR128_4 temperature: -159
2017-12-09_13:33:25 THR128_4 battery: low
2017-12-09_13:33:25 THR128_4 T: -159 BAT: low
2017-12-09_13:33:25 THR128_4 RSSI: -81.5
2017-12-09_13:33:25 THR128_4 DMSG: 500A4D400BA4F808004D00
2017-12-09_13:33:25 THR128_4 RAWMSG: Mc;LL=-2701;LH=3157;SL=-1254;SH=1667;D=D351FF6A;C=1463;L=32;R=241;
2017-12-09_13:33:55 THR128_4 temperature: -159
2017-12-09_13:33:55 THR128_4 battery: low
2017-12-09_13:33:55 THR128_4 T: -159 BAT: low
2017-12-09_13:33:55 THR128_4 RSSI: -79.5
2017-12-09_13:33:55 THR128_4 RAWMSG: Mc;LL=-2596;LH=3210;SL=-1246;SH=1677;D=D351FF6A;C=1454;L=32;R=245;
2017-12-09_13:33:55 THR128_4 DMSG: 500A4D400BA4F808004D00
2017-12-09_13:34:25 THR128_4 temperature: -158.9
2017-12-09_13:34:25 THR128_4 battery: low
2017-12-09_13:34:25 THR128_4 T: -158.9 BAT: low
2017-12-09_13:34:25 THR128_4 RSSI: -82
2017-12-09_13:34:25 THR128_4 RAWMSG: Mc;LL=-2614;LH=3189;SL=-1256;SH=1674;D=D391FFAA;C=1455;L=32;R=240;
2017-12-09_13:34:25 THR128_4 DMSG: 500A4D400B94F808004C00
2017-12-09_13:34:26 THR128_4 temperature: -158.9
2017-12-09_13:34:26 THR128_4 battery: low
2017-12-09_13:34:26 THR128_4 T: -158.9 BAT: low
2017-12-09_13:34:26 THR128_4 RSSI: -78.5
2017-12-09_13:34:26 THR128_4 DMSG: 500A4D400B94F808004C00
2017-12-09_13:34:26 THR128_4 RAWMSG: Mc;LL=-2637;LH=3186;SL=-1241;SH=1688;D=D391FFAA;C=1458;L=32;R=247;

Er scheint nun wo anderes noch was zu verschieben. Habe ich noch nicht genauer geschaut.
Würden dir bzw. EUCH MU Nachrichten Erkenntnisse nützen?

Verbos 5 - Ausgabe
>2017.12.09 13:38:55 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2715;LH=3161;SL=-1241;SH=1681;D=B07800D4;C=1466;L=30;R=208;
2017.12.09 13:38:55 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1466 RSSI -98 -> OSV1
2017.12.09 13:38:55 5: Arduino_Nano_433Mhz: extracted data 10110000011110000000000011010100 (bin)
2017.12.09 13:38:55 5: Arduino_Nano_433Mhz: protocol does not match return from method: ( message is to short)
2017.12.09 13:38:55 4: Arduino_Nano_433Mhz/msg READ: DMc;P0=-2630;P3=1677;P4=-1259;P7=3191;D=563470743034343434743434303434343434343434343474307071
2017.12.09 13:38:55 5: Arduino_Nano_433Mhz/noMsg Parse: DMc;P0=-2630;P3=1677;P4=-1259;P7=3191;D=563470743034343434743434303434343434343434343474307071
2017.12.09 13:38:55 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2630;LH=3191;SL=-1259;SH=1677;D=D3E1FFCA;C=1459;L=32;R=245;
2017.12.09 13:38:55 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1459 RSSI -79.5 -> OSV1
2017.12.09 13:38:55 5: Arduino_Nano_433Mhz: extracted data 11010011111000011111111111001010 (bin)
2017.12.09 13:38:55 4: Arduino_Nano_433Mhz: OSV1 input data: 11010011111000011111111111001010
2017.12.09 13:38:55 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.09 13:38:55 4: Arduino_Nano_433Mhz: converted to hex: (500A4D400B74F808004A00)
2017.12.09 13:38:55 5: Arduino_Nano_433Mhz Dispatch: 500A4D400B74F808004A00, test ungleich: disabled
2017.12.09 13:38:55 5: Arduino_Nano_433Mhz Dispatch: 500A4D400B74F808004A00, -79.5 dB, dispatch
2017.12.09 13:38:55 5: Arduino_Nano_433Mhz: dispatch 500A4D400B74F808004A00
2017.12.09 13:38:55 5: OREGON: decoding delay=30 hex=500A4D400B74F808004A00
2017.12.09 13:38:55 5: OREGON: sensor_id=0a4d BitsMsg=80 Bits=80
2017.12.09 13:38:55 5: OREGON: checksum2 = 74 berechnet: 74
2017.12.09 13:38:55 4: THR128_4 decoded Oregon: T: -158.7 BAT: low

Ralf9, 2017-12-09:
ich denke wir benötigen keine weiteren Erkenntnisse

sidey79, 2017-12-09:
Ich nehme an, wir müssten das Signal in FHEM invertieren.

Vielleicht kann man es auch daran festmachen, ob es mit 00 oder 11 beginnt?

sidey79, 2017-12-09:
@homeautouser

Habe die Logs erst gerade gesehen. Das Problem ist die Polarität. Das war mir bewusst.
Ich weiss nur noch nicht, an welcher Stelle man hier etwas ändert. Firmware oder in FHEM..

HomeAutoUser, 2017-12-15:
Hallo @sidey79 / @Ralf9 ,
ich habe nun den [neuen Commit](https://github.com/RFD-FHEM/SIGNALDuino/tree/1141fe7299a040bcd98f7003f56b3e59c673fa03) von @sidey79 ausprobiert und lasse ihn laufen.
Da die Erkennung überarbeitet wurde, so ist mir immer noch aufgefallen, das bezogen auf den THR 128 ständig nach einer Nachricht auch parallel die "invertierte Version" des Sensors immer noch empfangen wird. Ich hatte damals wie @Ralf9 beschrieben hatte, den Workaround gemacht, diesen Sensor auch als invertiert in die Signalduino.pm als Protocoll zu integrieren. Mit der neuen Firmware empfange ich nun jedes mal den Sensor "normal" und "zeitgleich auch die inertierte Variante". Ich vermute, die Invertierung ist eine Widerholung, welche invertiert interprätiert wird.

normale
>2017-12-15_20:24:28 THR128_1 T: 6.8 BAT: ok
2017-12-15_20:24:28 THR128_1 DMSG: 500A4D1004800600002400
2017-12-15_20:24:28 THR128_1 RAWMSG: MC;LL=-2709;LH=3164;SL=-1227;SH=1692;D=D3E9FFC6;C=1465;L=32;R=246;
2017-12-15_20:24:28 THR128_1 RSSI: -79

invertierte
>2017-12-15_20:24:28 THR128_4 temperature: -159.7
2017-12-15_20:24:28 THR128_4 battery: low
2017-12-15_20:24:28 THR128_4 T: -159.7 BAT: low
2017-12-15_20:24:28 THR128_4 LastActionTime: 2017-12-15 20:24:28
2017-12-15_20:24:28 THR128_4 RAWMSG: MC;LL=-2709;LH=3164;SL=-1227;SH=1692;D=D3E9FFC6;C=1465;L=32;R=246;
2017-12-15_20:24:28 THR128_4 RSSI: -79
2017-12-15_20:24:28 THR128_4 DMSG: 500A4D400B74F908004B00

Ralf9, 2017-12-15:
Wenn alle Nachrichten normal (ID 18) mit plausiblen Temperaturen empfangen werden und mit dem Workaround (ID 18.1) unplausible Temperaturen empfangen werden, dann wird der Workaround mit der ID 18.1 nicht mehr benötigt.

sidey79, 2017-12-15:
@HomeAutoUser 
Danke für den Hinweis.  Ralf und ich hatten uns jetzt erst mal drauf konzentriert, alle Bits vom Beginn einer Nachricht zu erkennen. An Wiederholungen und was das Bedeuten könnte, haben zumindest ich noch nicht bei dieser Anpassung gedacht.

Bei dem V1 Protokoll wird die Wiederholung doch Inklusive Preamble gemacht oder?

HomeAutoUser, 2017-12-15:
@sidey79,
>Bei dem V1 Protokoll wird die Wiederholung doch Inklusive Preamble gemacht oder?

Ich denke ja.
[Laut Protocol](http://www.osengr.org/WxShield/Downloads/Weather-Sensor-RF-Protocols.pdf) ja aber mit "Einschränkung vermutlich?" 
1) "The preamble contains twelve “1” bits. "
2) "Message Repeated: Yes"
3) "These sensors also repeat each message once, but do not repeat each bit. "

2017.12.16 00:31:58 4: Nano_433Mhz/msg READ: MC;LL=-2739;LH=3130;SL=-1260;SH=1687;D=DFD1FFE1;C=1469;L=32;R=5;
2017.12.16 00:31:58 4: Nano_433Mhz: Found manchester Protocol id 18 clock 1469 RSSI -71.5 -> OSV1
2017.12.16 00:31:58 5: Nano_433Mhz: extracted data 00100000001011100000000000011110 (bin)
2017.12.16 00:31:58 4: Nano_433Mhz: OSV1 input data: 00100000001011100000000000011110
2017.12.16 00:31:58 4: Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.16 00:31:58 4: Nano_433Mhz: converted to hex: (500A4D1004400700002100)
2017.12.16 00:31:58 5: Nano_433Mhz Dispatch: 500A4D1004400700002100, test ungleich: disabled
2017.12.16 00:31:58 5: Nano_433Mhz Dispatch: 500A4D1004400700002100, -71.5 dB, dispatch
2017.12.16 00:31:58 5: Nano_433Mhz: dispatch 500A4D1004400700002100
2017.12.16 00:31:58 4: Nano_433Mhz: Found manchester Protocol id 18.1 clock 1469 RSSI -71.5 -> OSV1
2017.12.16 00:31:58 5: Nano_433Mhz: extracted data 11011111110100011111111111100001 (bin)
2017.12.16 00:31:58 4: Nano_433Mhz: OSV1 input data: 11011111110100011111111111100001
2017.12.16 00:31:58 4: Nano_433Mhz: OSV1 protocol id (18.1) translated to RFXSensor format
2017.12.16 00:31:58 4: Nano_433Mhz: converted to hex: (500A4D400BB4F808004E00)
2017.12.16 00:31:58 5: Nano_433Mhz Dispatch: 500A4D400BB4F808004E00, test ungleich: disabled
2017.12.16 00:31:58 5: Nano_433Mhz Dispatch: 500A4D400BB4F808004E00, -71.5 dB, dispatch
2017.12.16 00:31:58 5: Nano_433Mhz: dispatch 500A4D400BB4F808004E00
2017.12.16 00:31:58 4: Nano_433Mhz/msg READ: MC;LL=-2730;LH=3153;SL=-1240;SH=1677;D=BFA3C;C=1466;L=18;R=3;

sidey79, 2017-12-16:
@HomeAutoUser 

Keine Ahnung, wieso jemand behauptet, dass die Preamble aus "1" Bits bestehen soll.
Wenn das so wäre, dann hätte derjenige der sich das überlegt hat, die Daten doch bestimmt mit der gleichen Taktrate gesendet wie das übliche Signa. Ist aber auch egal.

Bei mir klappt es problemlos, wenn ich Folgendes in den Puffer schiebe 
`MU;P0=-27224;P1=1673;P2=-1260;P3=-4304;P4=5712;P5=-6752;P6=3145;P7=-2718;D=12121212121212121212121345126712671212121267621712121212121212121212621712126;\n`

Kommt immer:
`MC;LL=-2718;LH=3145;SL=-1260;SH=1673;D=DBE9FFCE;C=1465;L=32;R=0`

als Ergebnis.

sidey79, 2017-12-17:
Also ich habe das Problem hier bei mir scheinbar auch, dass bei zwei Übertragungen eine invertiert ist.

```
2017.12.17 12:10:08.364 4: RTHN318_13_5 decoded Oregon: T: 18.5 BAT: ok
2017.12.17 12:10:08.327 4: sduino: OSV2 protocol converted to hex: (449ADC53135018C0433) with length (76) bits
2017.12.17 12:10:08.311 4: sduino: OSV2 protocol detected: preamble_pos = 34
2017.12.17 12:10:08.284 4: sduino: Found manchester Protocol id 10 clock 489 RSSI -72 -> OSV2o3
2017.12.17 12:10:08.267 4: sduino/msg READ: MC;LL=-1012;LH=941;SL=-525;SH=457;D=AAAAAAAA99A596A6A9666965556655A55556A956694;C=489;L=170;R=4;
2017.12.17 12:10:08.019 4: sduino: Found manchester Protocol id 58 clock 488 RSSI -72 -> tfa 30.3208.0 
2017.12.17 12:10:07.979 4: sduino: Found manchester Protocol id 52 clock 488 RSSI -72 -> OS_PIR
2017.12.17 12:10:07.939 4: sduino: Found manchester Protocol id 12.1 clock 488 RSSI -72 -> Hideki protocol not invert
2017.12.17 12:10:07.899 4: sduino: Found manchester Protocol id 12 clock 488 RSSI -72 -> Hideki protocol
2017.12.17 12:10:07.884 4: OREGON: ERROR: Unknown sensor_id=6b72 bits=68 message='446B724F4D4061000FD'.
2017.12.17 12:10:07.849 4: sduino: OSV2 protocol converted to hex: (446B724F4D4061000FD) with length (76) bits
2017.12.17 12:10:07.833 4: sduino: OSV2 protocol detected: preamble_pos = 29
2017.12.17 12:10:07.806 4: sduino: Found manchester Protocol id 10 clock 488 RSSI -72 -> OSV2o3
2017.12.17 12:10:07.790 4: sduino/msg READ: MC;LL=-1024;LH=937;SL=-535;SH=436;D=555555559969A5655A665A6AAA66A96AAAA55AA65A;C=488;L=168;R=4;

```

Ralf9, 2017-12-17:
Wie sehen bei deaktiviertem MC-Decder die MU-Nachrichten der beiden Übertagungen (invertiert + nicht invertiert) aus?

sidey79, 2017-12-18:
Ich habe mir jetzt zwei Signalduinos an einen Empfänger gemacht. Jetzt lasse ich bei einem MC aktiv und bei dem 2. MC inaktiv.

Die Baudrate habe ich mir mal wieder auf 500000 hochgestellt. Da geht sonst zu viel verloren.

sidey79, 2017-12-18:
Habe mir jetzt mal das mit der Wiederholung angesehen.

Im Puffer finde ich das, für die Wiederholung:

`MU;P0=-5848;P1=420;P2=-1040;P3=897;53232323232323232323232323232323412143412143412143232341232143234123214341`

@Ralf9 
Ich glaube das wir die lange Pause hier falsch auswerten. :(

sidey79, 2017-12-19:
So, ich mach jetzt debugging manuell:

Habe folgendes von einem OSV2 Sensor im Puffer. Signal stimmt, da ich zwei identische Übertragungen habe. (Es passt nur eine komplett in den Puffer!)

```
	Signalpuffer:
		P0: 2*[178] P1: 1*[-112] P2: 87*[-1034] P3: 87*[922] P4: 38*[-547] P5: 38*[425] P6: 1*[-9356]	
	23232323232323234525434525434525432323452325432345232543452323232325434523232323232323232543452323232543452543234523254323452325434523232323232543452325432323452323232323232325432363232323232323232323232323232323452543452543452543232345232543234523254

```

Manuell Demoduliert kommen folgende Bits heraus (. Bedeutet, dass diese Signalfolge kein Bit demodulier):
`	0101010101010101.10.01.10.01.10.010101.1010.0101.1010.01.1010101010.01.101010101010101010.01.10101010.01.10.0101.1010.0101.1010.01.101010101010.01.1010.010101.1010101010101010.0101
`

Der Signalduino demoduliert wie folgt:
`MC;LL=-1034;LH=922;SL=-547;SH=425;D=555533352D2CAACAAAACAB34B4B2AACB52AAAD0;C=487;L=153;R=0;
`
Wenn ich das in Bits umwandele kommt folgendes Heraus:
`	0101010101010101 00 11 00 11 00 110101 001011010 0101 10 0101010101 10 010101010101010101 10010101011 00 1101001 0110 1001 0110 01 010101010110 01 0110 101001 010101010101011010000
`
Es reichen die ersten paar Bits und schon sehen wir, irgendwas stimmt nicht:

```
0101010101010101.10.01 
0101010101010101 00 11
```

Ralf9, 2017-12-20:
Das einzige was an der OSV2 Nachricht nicht passt, ist daß am Anfang ein Teil der Preamble fehlt. Ich habe die Preamble von der Wiederholung genommen, dann passt es.
Ich habe die Nachricht dann mit "get sduino raw SR;..." von einem sduino zu einem andern gesendet

```
SR;P2=-1034;P3=922;P4=-547;P5=425;P6=-9356;D=32323232323232323232323232323234525434525434525432323452325432345232543452323232325434523232323232323232543452323232543452543234523254323452325434523232323232543452325432323452323232323232325432363232;
```

Bei dem anderen sduino habe ich dann folgendes empfangen:
```
2017.12.20 01:28:49.254 4 : sduino/msg Read: MC;LL=-1041;LH=908;SL=-559;SH=399;D=AAAAAAAA666A5A5955955559566969655596A5555A;C=484;L=167;
2017.12.20 01:28:49.254 4 : sduino: Found manchester Protocol id 10 clock 484 -> OSV2o3
2017.12.20 01:28:49.255 4 : sduino: OSV2 protocol detected: preamble_pos = 32
2017.12.20 01:28:49.255 4 : sduino: OSV2 protocol converted to hex: (44EA4C104068269003C) with length (76) bits
2017.12.20 01:28:49.255 5 : OREGON: decoding delay=342 hex=44EA4C104068269003C
2017.12.20 01:28:49.255 5 : OREGON: sensor_id=ea4c BitsMsg=68 Bits=64
2017.12.20 01:28:49.255 5 : OREGON: checksum1 = 57 berechnet: 57
2017.12.20 01:28:49.255 4 : THN132N_1 decoded Oregon: T: 26.6 BAT: ok
```

sidey79, 2017-12-20:
Ich verstehe was Du gemacht hast, aber das Problem liegt an einer anderen Stelle.

Ich habe mir das gestern noch angesehen.
Je nachdem, wie die Preamble empfangen wird, kommt es zu einer Demodulierung die nicht passt.

1. Musst vor dem ersten Manchester Signalpuls noch etwas Zusätzlich im Puffer sein
2. Je nachdem der erste Manchester Puls ein Low oder high ist, passt dann das ganze Signal nicht.

Dass ein Teil der Preamble fehlt ist halt eine Gegebenheit.

Problematisch ist halt, dass man bei der Wiederholung einen langen Low  Pegel als 1. im Puffer hat, der dann von einem langen high Pegel gefolgt wird. Dadurch ist das 1. Bit eine 1

Worüber ich noch unsicher bin ist, ob diese Moduliert überhaupt richtig ist.
Wäre der 1. Low Pegel nicht da, würde das 1. Bit als 0 erkannt.

sidey79, 2017-12-20:
Woran ich im Moment noch überlege ist, ob man die Sendepause zwischen den Wiederholungen in irgend einer besonderen Art berücksichtigt.

sidey79, 2017-12-21:
@HomeAutoUser 

So, wie sieht es denn mit dem aktuellen commit aus? 
7f34f53f96fd401392f2a54cbf1abf837839f5d4

HomeAutoUser, 2017-12-23:
Hallo @sidey79,
> So, wie sieht es denn mit dem aktuellen commit aus?

Das Verhalten ist bisher unverändert. Es trifft eine Message ein und Parallel dazu auch eine invertierte Message.

sidey79, 2017-12-23:
Das habe ich erwartet.
Eine von beiden wird aber richtig dekoriert oder?

HomeAutoUser, 2017-12-23:
> Eine von beiden wird aber richtig dekoriert oder?
Genau richtig ;-)

sidey79, 2017-12-26:
@HomeAutoUser 
Ich habe die DMc Ausgabe noch mal in der FW aktiviert. Kannst Du es damit noch mal laufen lassen.
Ich komme irgendwie nicht dahinter, wie das Signal ankommt, dass es zu einer invertierten Ausgabe kommt.

HomeAutoUser, 2017-12-28:
Hallo,
nachdem ich nun mein System erstmal wieder neu herstellen durfte bin ich wieder arbeitsbereit ;-)

@sidey79 hier hast du jeweils ein Empfang
**VERBOSE 4**
>2017.12.28 19:59:32 4: Arduino_Nano_433Mhz/msg READ: DMC;P2=-1252;P3=1683;P4=-4302;D=1234323232323232323234563270723032723032723230323232323232323272323032707
2017.12.28 19:59:32 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2716;LH=3150;SL=-1252;SH=1683;D=D331FF1A;C=1466;L=32;R=244;
2017.12.28 19:59:32 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1466 RSSI -80 -> OSV1
2017.12.28 19:59:32 4: Arduino_Nano_433Mhz: OSV1 input data: 11010011001100011111111100011010
2017.12.28 19:59:32 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.28 19:59:32 4: Arduino_Nano_433Mhz: converted to hex: (500A4D400BC4F808004F00)
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18.1 clock 1466 RSSI -80 -> OSV1
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: OSV1 input data: 00101100110011100000000011100101
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: OSV1 protocol id (18.1) translated to RFXSensor format
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: converted to hex: (500A4D1004300700002000)
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz/msg READ: DMC;P0=-1242;P1=1692;P5=3196;P6=-2621;D=012101010101012341056501610501610501016101010101010101050101610565
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2621;LH=3196;SL=-1242;SH=1692;D=D331FF1A;C=1458;L=32;R=243;
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1458 RSSI -80.5 -> OSV1
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: OSV1 input data: 11010011001100011111111100011010
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: converted to hex: (500A4D400BC4F808004F00)
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18.1 clock 1458 RSSI -80.5 -> OSV1
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: OSV1 input data: 00101100110011100000000011100101
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: OSV1 protocol id (18.1) translated to RFXSensor format
2017.12.28 19:59:33 4: Arduino_Nano_433Mhz: converted to hex: (500A4D1004300700002000)

>2017.12.28 20:00:02 4: Arduino_Nano_433Mhz/msg READ: DMC;P0=3145;P1=-2727;P3=-1250;P4=1675;D=23434343434345674301034143434301034341434343434343434343434301010
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2727;LH=3145;SL=-1250;SH=1675;D=D3D1FFEA;C=1466;L=32;R=243;
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1466 RSSI -80.5 -> OSV1
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: OSV1 input data: 11010011110100011111111111101010
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: converted to hex: (500A4D400BB4F808004E00)
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18.1 clock 1466 RSSI -80.5 -> OSV1
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: OSV1 input data: 00101100001011100000000000010101
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: OSV1 protocol id (18.1) translated to RFXSensor format
2017.12.28 20:00:02 4: Arduino_Nano_433Mhz: converted to hex: (500A4D1004400700002100)

**Ausgabe bei VERBOSE 5**
>2017.12.28 20:04:33 4: Arduino_Nano_433Mhz/msg READ: DMC;P1=1678;P2=-1268;D=0121212121212121212121213451267126712621712621217121212121212121262121767126
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz/noMsg Parse: DMC;P1=1678;P2=-1268;D=0121212121212121212121213451267126712621712621217121212121212121262121767126
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-2715;LH=3138;SL=-1268;SH=1678;D=DB31FF16;C=1466;L=32;R=244;
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18 clock 1466 RSSI -80 -> OSV1
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz: extracted data 11011011001100011111111100010110 (bin)
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: OSV1 input data: 11011011001100011111111100010110
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: OSV1 protocol id (18) translated to RFXSensor format
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: converted to hex: (500A4D400BC4F808004F00)
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz Dispatch: 500A4D400BC4F808004F00, test ungleich: disabled
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz Dispatch: 500A4D400BC4F808004F00, -80 dB, dispatch
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz: dispatch 500A4D400BC4F808004F00
2017.12.28 20:04:33 5: OREGON: decoding delay=60 hex=500A4D400BC4F808004F00
2017.12.28 20:04:33 5: OREGON: sensor_id=0a4d BitsMsg=80 Bits=80
2017.12.28 20:04:33 5: OREGON: checksum2 = 79 berechnet: 79
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18.1 clock 1466 RSSI -80 -> OSV1
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz: extracted data 00100100110011100000000011101001 (bin)
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: OSV1 input data: 00100100110011100000000011101001
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: OSV1 protocol id (18.1) translated to RFXSensor format
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: converted to hex: (500A4D1004300700002000)
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz Dispatch: 500A4D1004300700002000, test ungleich: disabled
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz Dispatch: 500A4D1004300700002000, -80 dB, dispatch
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz: dispatch 500A4D1004300700002000
[... cut ...]
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz: protocol does not match return from method: ( message is to short)
2017.12.28 20:04:33 4: Arduino_Nano_433Mhz: Found manchester Protocol id 18.1 clock 1466 RSSI -80.5 -> OSV1
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz: extracted data 01101100110001111111110001011011 (bin)
2017.12.28 20:04:33 5: Arduino_Nano_433Mhz: protocol does not match return from method: ( message is to short)

Solltest du keine Hinweise bzw Erkenntnisse finden. So könnte ich mit @elektron-bbs Mitte Januar mal das Ozi dran hängen. ;-)

elektron-bbs, 2017-12-28:
Da kommen aber jeweils 2 identische Nachrichten vom SIGNALduino. Je nachdem, welche die richtige ist, solltest du mal Protokolldefinition 18 oder 18.1 deaktivieren.

sidey79, 2017-12-28:
@HomeAutoUser  @elektron-bbs 

Sehe ich auch so. Die 1. Nachricht sowie die Wiederholung  werden identisch Empfangen und an FHEM übergeben. Das ist super. So soll es sein.

Die Protokoll ID 18.1 würde ich mal schätzen führt auch zu den richtigen Temperaturen. Die ID 18 zu den falschen.

HomeAutoUser, 2017-12-29:
Kann es sein @sidey79, das durch die Arbeiten am Decoder sich bei der Erkennung die "Polarität" geändert hat? Ich hatte die Protokolldefinition 18.1 bei mir entfernt. Da kam der Sensor auf einmal nicht mehr, da der "invertierte durchkam". Nachdem ich die Protokolldefinition 18 nun auskommentierte und die 18.1 am laufen habe, so kam dieser sofort mit richtigen Werten wieder.

>"18"    => 			## Oregon Scientific v1
		# {
            # name			=> 'OSV1',	
			# id          	=> '18',
			# clockrange     	=> [1400,1500],			# min , max
			# format 			=> 'manchester',	    # tristate can't be migrated from bin into hex!
			# preamble		=> '',					
			# clientmodule    => 'OREGON',
			# modulematch     => '^[0-9A-F].*',
			# length_min      => '31',
			# length_max      => '32',
			# polarity        => 'invert',		    # invert bits
			# method          => \&SIGNALduino_OSV1   # Call to process this message
		# },
	"18.1"    => 			## Oregon Scientific v1
		{
            name			=> 'OSV1',	
			id          	=> '18',
			clockrange     	=> [1400,1500],			# min , max
			format 			=> 'manchester',	    # tristate can't be migrated from bin into hex!
			preamble		=> '',					
			clientmodule    => 'OREGON',
			modulematch     => '^[0-9A-F].*',
			length_min      => '31',
			length_max      => '32',
			#polarity        => 'invert',		    # invert bits
			method          => \&SIGNALduino_OSV1   # Call to process this message
		},

Da von Anfang an die Definition 18 mit "polarity        => 'invert'," versehen war, so wäre ja nun die 18.1 die nicht invertierte. Diese Erkenntnis müsste ggf mal gegengeprüft werden um die SIGNALduino.pm anzupassen.

HomeAutoUser, 2017-12-29:
Ich habe mich nun wieder mit der Thematik beschäftigt, weil erneut Ausreißer mit Diff 5 Grad erschießnen sind. Da kam mir die Idee, LongID mal zu aktivieren um das ganze genauer unter die Lupe zu nehmen.

LongID definiert (im SIGNALduino Device)
>longids | 41_OREGON

aber ich erhalte noch keine longids. Der Sensor wird weiterhin mit der Kanalnummer angelegt. Im Modul selbst ist auch die Funktion drin auf den ersten Blick. Habe ich einen Denkfehler? Bei den Modulsen SD_WS unsw. funktionierte dies dort an der Stelle.

elektron-bbs, 2017-12-29:
Ich denke, du musst THR128 und nicht OREGON im Attribut "longids" verwenden.
# Device Namen sehen z.B. so aus: SD_WS07_TH_3 for channel 3.
 attr cul longids SD_WS07

HomeAutoUser, 2017-12-29:
Ich habe alle Varianten ausprobiert! Auch
>THR128_1

oder ...
>OREGON_THR_128_1

absolut nicht! Ich vermute da liegt ein anderes Problem vor.
attr SIGNALduino longids ...

HomeAutoUser, 2017-12-29:
Ich kann auf jedenfall feststellen, wenn die orginal Definition des Protokolles 18
>	"18"    => 			## Oregon Scientific v1
		{
            name			=> 'OSV1',	
			id          	=> '18',
			clockrange     	=> [1400,1500],			# min , max
			format 			=> 'manchester',	    # tristate can't be migrated from bin into hex!
			preamble		=> '',					
			clientmodule    => 'OREGON',
			modulematch     => '^[0-9A-F].*',
			length_min      => '31',
			length_max      => '32',
			polarity        => 'invert',		    # invert bits
			method          => \&SIGNALduino_OSV1   # Call to process this message
		},

genutzt wird, so kommt ein Sensor mit falschen Werten heraus.
>THR128_0b_4 | T: -159.7 BAT: low

Sobald ich in der Definition die Zeile
>#polarity        => 'invert',		    # invert bits

ändere und somit die Definition die Polarität "normal" nutzt, so kommt der richtige Sensor heraus.

> THR128_04_1 | T: 7.7 BAT: ok

@sidey79, nun die Frage an dich. Kann es sein, das im laufe des Decoderumbaus, sich dort bei der Erkennung die Polarität geändert hat. Das ganze Verhalten müsste auch mal durchspielt werden mit einem anderen OREGON Sensor. @Ralf9 , besitzt du einen? Wir müssen herausfiltern, ob das Verhalten nur bei diesem Sensor so ist.

Was die "Empfangssprünge" von Werten angeht, das muss ich auf längere Zeit beobachten.

[87 more replies]

commit 9f4a8218 (2018-02-11): ManchesterpatternDecoder

doDecode() modified preamble detection again, to better support all variants from #82

Try to fix #75
closed 2019-12-03 by sidey79