RFD-FHEM/SIGNALDuino #76: FS20|FHT|FHTTK - Erkennung nur bei deaktiven MC?

HomeAutoUser, 2017-11-15:
Hallo, @sidey79 
1) ich habe mal deine [neue FW getestet](https://github.com/RFD-FHEM/SIGNALDuino/tree/8126d15fefb4e8db28d48990ef549b425b3429f4) und diese auf dem 868Mhz Empfänger am laufen.
Ist es normal bzw. gewollt, das ich die Nachrichten nur komprimiert sehe? Kann man dies abschalten? Eine Überprüfung ist somit nicht mehr möglich.

**- alle Decoder aktiv -**
>2017.11.15 22:09:22 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;�Ӂ;���;���;d!!!!!!!!!!!#A#CA!#CA!#CCA!#A#CCCA#CCA!#A!!!!!!#CA!#CA#A!#A;C1;RF;
2017.11.15 22:09:22 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;��;���;���;d!!!!!!!!!!!#A#CA!#CA!#CCA!#A#CCCA#CCA!#CA!!!!!#A#A#CA#A!!!;C1;RA;
2017.11.15 22:09:22 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=35  seq_even[3]=43  seq_odd[3]=15MC;P1=-596;P3=-398;P4=396;P5=583;D=4;
2017.11.15 22:09:22 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-596;LH=583;SL=-398;SH=396;D=F55ABB7B6AAABFB7AB6;C=328;L=75;R=252;
2017.11.15 22:09:23 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=45  seq_even[2]=24  seq_odd[2]=-1Mu;���;���;�ς;�ۂ;d!!!!!!!!!!!$TTTQ$TTQ!$Q$Q!!$Q$Q$TTTTTTTTQ!!!!$Q!$Q!$TTQ$TQ;C1;R5;
2017.11.15 22:09:25 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;�Ѓ;��;��;�ڃ;d!!!!!!!!!#CA!!#A!!!#CA!#CCA!#CCA!!#CCCA#A!!!#A#CA#CA!!#CCCA#;C3;R11;
2017.11.15 22:10:06 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=381;P2=-413;P3=574;P4=-614;D=012121212121212121212121234343434123434343434341212341234123412121234343434341234121212121212341234343412341212121234;CP=1;R=48;
2017.11.15 22:10:06 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.15 22:10:06 4: Arduino_Nano_868Mhz: FHT80TF - door/window switch protolength 45
2017.11.15 22:10:06 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 70 dmsg TEF953E02 length 32 RSSI = -50
2017.11.15 22:10:06 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.15 22:10:06 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.15 22:10:06 1: PERL WARNING: Use of uninitialized value $returnvalues[0] in join or string at ./FHEM/00_SIGNALduino.pm line 4127.
2017.11.15 22:10:07 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=32  seq_even[2]=21  seq_odd[2]=56  seq_even[3]=25  seq_odd[3]=-1Mu;���;���;�ʂ;�ׂ;d###!!!!!!!!%eeea%eeeeea!%a%a%a!!%eeeea%ea!!!!%a!!%a%a!!!!!;C1;R32;
2017.11.15 22:10:18 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.11.15 22:10:38 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=34  seq_even[2]=43  seq_odd[2]=14  seq_even[3]=41  seq_odd[3]=56MC;P1=396;P4=-397;P5=602;P6=-620;D=17141414141414141414141414561414561414141414561414561456561414561414141414141414145614561414565614141414561456141414141414561;O;
2017.11.15 22:10:38 4: Arduino_Nano_868Mhz/msg READ: MC;LL=-620;LH=602;SL=-397;SH=396;D=F77EED77FEDD7DBFBAD;C=335;L=76;R=27;
2017.11.15 22:10:38 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=17  seq_odd[0]=71  seq_even[1]=14  seq_odd[1]=41  seq_even[2]=56  seq_odd[2]=45  seq_even[3]=-1  seq_odd[3]=61Mu;���;���;��;��;DVVVVVVVVVVVVVVVVV;C1;R1B;
2017.11.15 22:10:45 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=15  seq_even[3]=41  seq_odd[3]=-1  seq_even[0]=21  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=34  seq_even[2]=41  seq_odd[2]=-1Mu;���;���;��;�܂;d!!!!!!!!!!!#A!!#A#CCA!#A!!!!!#A!!!!!!!!#A#A!#CA!!!#A!#CCA!!#CA!!!!!Q!!!!!!!!!!!#A!!#A#CCA!#A!!!!!#A!!!!!!!!#A#A!#CA!!!#A!#CCA!;C1;RF8;O;
2017.11.15 22:10:46 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=45  seq_even[2]=24  seq_odd[2]=-1Mu;���;���;�Ђ;�܂;d!!!!!!!!!!!$Q!!$TQ!$TQ$Q$TQ$TTQ!$TQ!!!!!!!!!!$Q$TQ$TQ!$Q!!;C1;R5;
2017.11.15 22:10:46 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;���;�Ղ;�ׂ;d!!!!!!!!!!!#A!!#CA!#CA#A#CA#CCA!#CA!!!!#A!!!!#A!!!#CA!#A#A;C1;RFC;
2017.11.15 22:10:48 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=15  seq_even[3]=41  seq_odd[3]=-1  seq_even[0]=21  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=34  seq_even[2]=41  seq_odd[2]=-1Mu;���;���;��;��;d!!!!!!!!!!!#A!!#A#CA#A!!#A#CCA!!!!!!!!!#A#A!#CA!!!!!!!!!!#CA#CCCCCAQ!!!!!!!!!!!#A!!#A#CA#A!!#A#CCA!!!!!!!!!#A#A!#CA!!!!!!!!!!#;C1;RF9;O;
2017.11.15 22:10:48 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;���;���;��;d!!!!!!!!!!!#A!!#A#CA#A!!#A#CCA!#CCCCA#A!#CA!#CCCA!!#A!#A!!!#A#CCCCA;C1;RF9;
2017.11.15 22:10:48 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;���;��;��;d!!!!!!!!!!!#A!!#A#CA#A!!#A#CCA!#CCCCA#A!#CA!#CCCA!!#A!#A!!!#A#CCCCA;C1;RF8;
2017.11.15 22:10:48 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=45  seq_even[2]=24  seq_odd[2]=-1Mu;���;���;�݂;�;d!!!!!!!!!!!$Q!!$Q$TQ$Q!!$Q$TTQ!$TTTTQ$Q!$TTQ$TTQ!!!$Q!$Q!!!$TTTTTQ!;C1;R3A;
2017.11.15 22:10:48 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;���;��;�ւ;d!!!!!!!!!!!#A!!#A#CA#A!!#A#CCA!#A!!#A!!!#CA#A!#A!!!!!!!#CCCCA!#CCA!;C1;RF9;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-410;P1=574;P2=-605;P3=403;D=0123012121230301230303012303030301212121230301212303030303030301212121212123012121212;CP=3;R=57;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;���;��;�ق;d!!!!!!!!!!!#A!!#A#CA#A!!#A#CCA!#A!#A#CA!#CA!#CCCA!!#A!#A!#CCCCCA#CA;C1;RF9;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;���;��;��;d!!!!!!!!!!!#A!!#A#CA#A!!#A#CCA!#A!#A#CA!#CA!#CCCA!!#A!#A!#CCCCCA#CA;C1;RFD;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=56  seq_even[2]=25  seq_odd[2]=-1Mu;���;���;�؂;�͂;D!!!!!!!!!!!%a!!%a%ea%a!! ;C1;R3A;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-408;P1=584;P2=-592;P3=418;D=012303030123030303012121212303012123030303030303012121212123012121212;CP=3;R=58;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=32  seq_even[2]=41  seq_odd[2]=34  seq_even[3]=-1  seq_odd[3]=14Mu;��;���;���;�߂;D###A#AA#A###A#AAA##A##A#AA##AA##AAAA###A##A##AAAAAA#AA ;C2;RF9;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=45  seq_even[2]=24  seq_odd[2]=-1Mu;���;���;���;�Ȃ;d!!!!!!!!!!!$Q!!$Q$TQ$Q!!$Q$TTQ!$Q!$Q$TQ!$TTQ$TTQ!!!$Q!$Q!!!!!$TQ$TQ;C1;R3A;
2017.11.15 22:10:49 4: Arduino_Nano_868Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1Mu;���;���;��;���;d!!!!!!!!!!!#A!!#A#CA#A!!#CCA!#CCCCCA!!#CA!#CCCA!!#A!#A!!!#CA!!!!!;C1;RF9;
[... cut ...]
2017.11.15 22:28:00 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.11.15 22:28:00 4: Arduino_Nano_868Mhz: FHT80TF - door/window switch protolength 45
2017.11.15 22:28:00 4: Arduino_Nano_868Mhz: decoded matched MU Protocol id 70 dmsg T81460E82 length 32 RSSI = -78.5
2017.11.15 22:28:00 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2017.11.15 22:28:00 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate

sidey79, 2017-11-15:
@HomeAutoUser 

Hi, die Message Compression lässt sich über den Befehl CED deaktivieren und mit CER wieder aktivieren. Das ist schon länger so :)

HomeAutoUser, 2017-11-15:
@sidey79,
danke für den Hinweis mit der Message Compression ;-) Wir lernen nie aus.

Zwecks der MC Thematik wenn dieser aktiv ist und "die MU Nachrichten " modifiziert, so habe ich dir hier mal die reinen MU Nachrichten gesammelt. **Vielleicht lässt sich ja noch was finden wo der Decoder selber mitbekommt, HALT es ist kein MC.** Die reinen Daten sehen ja schonmal gut aus. Manchmal gibt es noch einen Overflow.

>2017.11.15 22:40:38 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=402;P2=-392;P3=591;P4=-598;D=012121212121212121212121234121212343412123434123412343412343434121234341212121212121212121212341234341234341212341212;CP=1;R=0;
2017.11.15 22:40:38 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=976;P1=-29176;P2=396;P3=-392;P4=605;P5=-589;D=012323232323232323232323234523232345452323454523452345452345454523234545232323232345232323232345232323234545232345234520;CP=2;R=2;
2017.11.15 22:40:45 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.11.15 22:41:25 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-29988;P1=396;P2=-405;P3=611;P4=-604;P5=-8940;D=0121212121212121212121212341212123412343434121234121212121212341212121212121212123412341212343412121212341212343412341212341234343434341512121212121212121212121234121212341234343412123412121212121234121212121212121212341234121234341212121234121234341234;CP=1;R=249;O;
2017.11.15 22:41:35 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=854;P2=-363;P3=355;P4=-861;D=0121212121212121212123434121212343412121234121212123434121212341234121234343412123434341212343412341234123434123434341234;CP=3;R=31;
2017.11.15 22:41:38 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-2172;P1=395;P2=-393;P3=589;P4=-600;D=012121212121212121212121234343434123434341212341234121212341234123434343434343434121212121212341234121212343434123412;CP=1;R=16;
2017.11.15 22:41:39 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-593;P1=381;P2=-2852;P3=180;P4=-394;P5=244;P7=601;D=1234545414141414147070707014707070141470147014141470147014707070707070707070141414141470141470141470707014707010;CP=1;R=19;
2017.11.15 22:41:45 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.11.15 22:41:50 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=402;P2=-386;P3=620;P4=-598;P5=-8936;D=0121212121212121212121212341212341212121212341212341234341212341212121212121212123412341212343412121212121234343412341212121234341212121512121212121212121212121234121234121212121234121234123434121234121212121212121212341234121234341212121212123434341234;CP=1;R=28;O;
2017.11.15 22:41:52 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-23104;P1=848;P2=-364;P3=371;P4=-857;D=0121212121212121212123434121212341212121234341212343434121234343412121234343434123412121212341234341234341212123434343412;CP=3;R=254;
2017.11.15 22:41:56 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-1053;P1=878;P4=-584;P5=400;P6=540;D=010101454505454101054145054101454501054;CP=5;R=254;
2017.11.15 22:41:56 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=520;P1=-565;P2=392;P3=-1070;P4=286;P5=882;D=012123212141214121512121212353212151232120;CP=2;R=213;
2017.11.15 22:42:24 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=404;P2=-10116;P3=-390;P4=612;P5=-614;D=0121313451345451345131313451345454513131313131313131313451345131345451313131313451345454513454545451345451313;CP=1;SP=2;R=242;m=2;
2017.11.15 22:42:24 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=404;P2=-10116;P3=-390;P4=612;P5=-614;D=12131313131313131313131313451313134513454513451313134513454545131313131313131313134513451313454513131313134513454545134545454513454513131;CP=1;R=242;
2017.11.15 22:42:25 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=431;P2=-351;P3=642;P4=-555;D=012121212121212121212121234123434121234341212343434121234123434343412343434121234121212121212123434121234341234121234;CP=1;R=13;
2017.11.15 22:42:26 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-22268;P1=429;P2=-359;P3=647;P4=-551;D=012121212121212121212121234123434121234341212343434121234123434343412343434121234341212121212123412341234341234121212;CP=1;R=13;
2017.11.15 22:42:38 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=397;P2=-391;P3=599;P4=-594;D=012121212121212121212121234343434123434343434341212341234123412121234343434341234121212121212341234343412341212121234;CP=1;R=45;
2017.11.15 22:42:39 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-17192;P1=404;P2=-394;P3=603;P4=-595;D=012121212121212121212121234343434123434343434341212341234123412121234343434341234341212121212341212123412341212121212;CP=1;R=45;
2017.11.15 22:42:42 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=394;P2=-407;P4=588;P5=-598;D=012121212121212121212121245451212124545451212454545121245121212454545454512124512121212121212451245124545454512124545;CP=1;R=248;
2017.11.15 22:42:43 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-8752;P1=392;P2=-394;P3=589;P4=-606;D=012121212121212121212121234341212123434341212343434121234121212343434343412123412341212121212341212343434343412123412;CP=1;R=248;
2017.11.15 22:42:45 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.11.15 22:43:20 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-31904;P1=400;P2=-394;P3=621;P4=-603;P5=-8984;D=0121212121212121212121212341212123412343434121234121212121212341212121212121212121212341212343412341212341212341234343412341234343412341512121212121212121212121234121212341234343412123412121212121234121212121212121212121234121234341234121234121234123434;CP=1;R=255;O;
2017.11.15 22:43:45 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
[... cut ...]
2017.11.15 22:44:43 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-415;P1=408;P2=-620;P5=585;P7=-10416;D=1212121012121212121212125212121252521212121212521212521210121212101212121252521212121212121212121212525212521010521017101010101010101010101010521010105252101010101052101052101010101010101010105252101010101010101010;CP=1;R=54;
2017.11.15 22:44:43 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-414;P1=389;P3=641;P4=-609;P6=-316;P7=-10488;D=0101034341034101634101710101010101010101010101034101010343410101010103410103410101010101010101010343410101010101010101010103434103410103410;CP=1;R=55;
2017.11.15 22:44:45 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0

HomeAutoUser, 2017-11-15:
Beispiel:
>2017.11.15 22:45:15 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=395;P2=-396;P3=595;P4=-624;P5=-8948;D=0121212121212121212121212341212123412343434121234121212121212341212121212121212123412341212343412121212341212341234341212341234343412121512121212121212121212121234121212341234343412123412121212121234121212121212121212341234121234341212121234121234123434;CP=1;R=253;O;

Hier ist die P5 die Trennung. Wenn ich es künstlich mal untereinander stelle sieht es wie folgt aus
>D=
0
121212121212121212121212341212123412343434121234121212121212341212121212121212123412341212343412121212341212341234341212341234343412121
5
12121212121212121212121234121212341234343412123412121212121234121212121212121212341234121234341212121234121234123434
;CP=1;R=253;O;

So wäre es mit vollständiger Trennung. Bitte nich irretieren lassen das der erste Teil länger ist. Laut Protokoll wird im Rest keine Checksumme übertragen.

sidey79, 2017-11-15:
@HomeAutoUser 

Ich weiss ja was Du willst, das mit dem Splitten der Nachrichten (aktuell MU) würde ich erst mal noch zurück stellen und mich erst mal um das MU / MC Thema kümmern.  Irgendwie bin ich noch am Überlegen ob ich diese FS20 Nachrichten vielleicht als MS übergebe...

Zu dem MC Decoder. Das Problem es, das Signal ist manchester Konform moduliert.
Was spricht denn dagegen, es manchester Kodiert zu verwerten? Das müsste doch gehen oder?

elektron-bbs, 2017-11-19:
In beiden letzten Firmware-Versionen (dev-r33_cc1101 und dev-r33-cc1101-mcfirtbitfix) scheint noch ein Bug zu sein. Mu-Nachrichten werden nicht dekomprimiert. Ich schätze, das liegt an den Debug-Ausgaben der Sequenzen vom MC-Dekoder. Wenn ich MC abschalte, funktioniert es.
[Bug_171119.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/1485453/Bug_171119.txt)

sidey79, 2017-11-19:
Ich vermute da fehlt ein Zeilenumbruch. Vermutlich wird der nur im Debug Modus ausgegeben und den habe ich wohl beendet (bis auf die Ausgabe der Sequenzen)

elektron-bbs, 2017-11-19:
Ich habe die Zeilen jetzt in die Debug-Ausgabe verfrachtet:

#if DEBUGDETECT >= 1
								DBG_PRINT("  seq_even["); DBG_PRINT(a); DBG_PRINT("]");
								DBG_PRINT("="); DBG_PRINT(sequence_even[a]);
								DBG_PRINT("  seq_odd["); DBG_PRINT(a); DBG_PRINT("]");
								DBG_PRINT("="); DBG_PRINT(sequence_odd[a]);
#endif

sidey79, 2017-11-25:
@elektron-bbs 

Ich nehme an, trotz deaktivierter Ausgabe, werden die Nachrichten immer noch als Manchester erkannt, wenn der MC Decoder aktiv ist. Richtig?

elektron-bbs, 2017-11-26:
Tja, das Verhalten ist unterschiedlich.
FHT wird bei mir eigentlich zuverlässig dekodiert.
[FHT_fhem_2017-11-26.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1503864/FHT_fhem_2017-11-26.log)
[FHT_5219-2017-11.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1503865/FHT_5219-2017-11.log)
Bei FS20 sieht es schlechter aus. Da kann er sich nicht entscheiden, mal MU, mal MC und MS.
[FS20_fhem_2017-11-26.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1503869/FS20_fhem_2017-11-26.log)
Das FS20-Protokoll gesendet von einem CUL wird gar nicht erkannt.
[CUL_FS20_fhem_2017-11-26.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1503875/CUL_FS20_fhem_2017-11-26.log)
Zum FHTTK kann ich nichts beitragen, da ich so etwas nicht habe. Dazu müsste sich @HomeAutoUser äußern.
Diese 3 Protokolle sind identisch, nur das bei FHTTK und FHT eine Wiederholung erfolgt und bei FS20 zwei.

elektron-bbs, 2017-11-26:
Mit der soeben veröffentlichten Firmware (commit c8fa0816f4db7ecc59038e8765c0de1a12a82ebd) werden Nachrichten vom FHT nicht mehr dekodiert. Die Puls-/Pausezeiten passen nicht mehr.
[FHT_fhem_2017-11-26_1.log](https://github.com/RFD-FHEM/SIGNALDuino/files/1503910/FHT_fhem_2017-11-26_1.log)

sidey79, 2017-11-26:
Mit dem letzten 
Commit kann das kaum Zusammenhängen.

Das Log kann ich auf dem Handy leider nicht öffnen. Ist Mal wieder irgendein Kompatibilitäts Probleme.

sidey79, 2017-11-26:
Entschuldige die Nachfrage, beim Compilieren hast Du aber ein 8 Mhz Modell ausgewählt?

elektron-bbs, 2017-11-27:
Definitiv ja, eben absichtlich mal die 16MHz ausgewählt - da geht gar nix:

2017.11.27 17:39:02 3: sduino868 reset
2017.11.27 17:39:02 3: Opening sduino868 device /dev/ttyUSB0
2017.11.27 17:39:02 3: Setting sduino868 serial parameters to 57600,8,N,1
2017.11.27 17:39:02 1: sduino868/define: /dev/ttyUSB0@57600
2017.11.27 17:39:02 1: sduino868/init: /dev/ttyUSB0@57600
2017.11.27 17:39:02 3: sduino868 device opened
2017.11.27 17:39:03 3: sduino868/init: disable receiver (XQ)
2017.11.27 17:39:04 3: sduino868/init: get version, retry = 0
2017.11.27 17:39:14 3: sduino868/init: get version, retry = 1
2017.11.27 17:39:24 3: sduino868/init: get version, retry = 2
2017.11.27 17:39:34 3: sduino868/init: get version, retry = 3
2017.11.27 17:39:34 2: sduino868/init retry count reached. Reset
2017.11.27 17:39:34 3: sduino868 reset

Kann eigentlich auch nicht, da die serielle ja dann mit doppelten Takt läuft.

Ich habe eben nochmal die jetzt aktuelle Version (Latest commit 736164f) probiert. Etwa eine halbe Stunde laufen lassen - keine Dekodierung vom FHT. Einmal kam eine Nachricht als MS, die passen könnte. Ein jetzt hier zur Verfügung stehender FHTTK wurde allerdings fehlerfrei empfangen.
Jetzt wieder unverändert die alte Firmware drauf... und siehe da, es geht wieder:

2017-11-27_17:12:09 FHT_5219 actuator: 13%
2017-11-27_17:12:09 FHT_5219 DMSG: 810c04xx0909a00152190000B621
2017-11-27_17:12:09 FHT_5219 RSSI: -72.5
2017-11-27_17:14:04 FHT_5219 actuator: 13%
2017-11-27_17:14:04 FHT_5219 RSSI: -58.5
2017-11-27_17:14:04 FHT_5219 DMSG: 810c04xx0909a00152190000B621
2017-11-27_17:44:53 FHT_5219 actuator: 13%
2017-11-27_17:44:53 FHT_5219 RSSI: -58.5
2017-11-27_17:44:53 FHT_5219 DMSG: 810c04xx0909a00152190000B620
2017-11-27_17:46:48 FHT_5219 actuator: 13%
2017-11-27_17:46:48 FHT_5219 RSSI: -58.5
2017-11-27_17:46:48 FHT_5219 DMSG: 810c04xx0909a00152190000B620

HomeAutoUser, 2017-11-27:
Hallo,

hier ist mal wieder ein klassischer Fall MS / MU Erkennung bei einem FS20 Protocoll bei deaktiven MC Decoder.

>2017.11.27 19:15:25 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=391;P2=-403;P4=582;P5=-610;P6=-9138;D=21612121212121212121212121245124545121245451212451212454512124512121212121212121212121212121212121212121212121245124512;CP=1;SP=6;R=252;O;
2017.11.27 19:15:25 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-401;P1=572;P2=-605;P3=386;D=012301212303012123030123030121230301230303030303030303030303030303030303030303030303012301230;CP=3;R=246;
2017.11.27 19:15:25 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.27 19:15:25 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=401;P2=-425;P4=-329;P5=603;P6=-577;P7=-10400;D=21712121212141212121412121256121412565612121212125612125612121212121212141212565614121212121212121212125656125612125612;CP=1;SP=7;R=43;O;
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-417;P1=418;P2=582;P3=-588;P4=322;D=010101023101040232310104010102310102310101040101010401010232310401010104010101010102323102310102310;CP=1;R=43;
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=407;P2=-402;P4=603;P5=-577;P7=-10412;D=21712121212121212121212121245121212454512121212124512124512121212121212121212454512121212121212121212124545124512124512;CP=1;SP=7;R=43;O;
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-400;P1=429;P2=593;P3=-614;P5=328;D=010101023101010232310101010102350102350101010101010101010232310101010101050101010102323102350102310;CP=1;R=43;
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=412;P2=-404;P5=588;P6=-596;P7=-10414;D=21712121212121212121212121256121212565612121212125612125612121212121212121212565612121212121212121212125656125612125612;CP=1;SP=7;R=42;O;
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-415;P1=404;P3=583;P4=-583;D=010101034101010343410101010103410103410101010101010101010343410101010101010101010103434103410103410;CP=1;R=42;
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.27 19:15:26 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=423;P2=-400;P3=328;P4=596;P5=-581;P7=-10412;D=21712323212123212121212121245121232454512121232124512124512121212121212121212454512321212123212121212324545124512124512;CP=1;SP=7;R=42;O;
2017.11.27 19:15:27 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-406;P1=326;P2=420;P3=596;P4=-581;D=010202034202020343420201020203410203420202010102020201020343420202020201020202020203434203420203420;CP=2;R=43;
2017.11.27 19:15:27 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2017.11.27 19:15:29 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0

sidey79, 2017-11-27:
Tia, mal ist der Syncpuls dabei und dann auch mal nicht... was könnte dazu denn führen?

sidey79, 2017-11-27:
@elektron-bbs 

Kannst Du noch mal genauer angeben, welche Version die ist, welche funktioniert?
Ist der Unterschied noch immer die Zeiten der Pulse?

elektron-bbs, 2017-11-27:
Das ist kein Syncpuls, sondern eine Pause bzw. Low-Pegel zwischen den Wiederholungen. Ein Syncpuls sollte am Beginn der Nachricht sein, oder liege ich da falsch? Bei den Beispielen fehlt offensichtlich der erste Protokollteil, der eigentlich meistens gekennzeichnet ist durch eine P0 von ca. 32000.

sidey79, 2017-11-27:
@HomeAutoUser 

Passt halt gut ins Schema eines Sync Pulses, da er vor einer Nachricht kommt.
P2=-400;P7=-10412;

`2017.11.27 19:15:26 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=423;P2=-400;P3=328;P4=596;P5=-581;P7=-10412;D=21712323212123212121212121245121232454512121232124512124512121212121212121212454512321212123212121212324545124512124512;CP=1;SP=7;R=42;O;
`

@elektron-bbs 
ich schau noch mal nach Änderungen seit dem 19.11

sidey79, 2017-11-27:
@elektron-bbs 

Ich habe nichts verändert, was Pulszeiten verändert.... außer gestern, da habe ich einen Werteüberlauf behoben :)

HomeAutoUser, 2017-11-27:
@sidey79 

>Passt halt gut ins Schema eines Sync Pulses, da er vor einer Nachricht kommt.

Was würdest du dann hier bei den Nachrichten als Sync definieren? Mein Versuch brachte bisher keine Erfolge mit   _sync => [-24.8,1],   # -10000,400_
Die Nachricht wird zwar als MS erkannt und weiterverabreitet die Routine aber bisher stimmen die CheckSummen noch nicht. Wenn ich das richtig überblicke, so ist es teilweise eine unterschiedliche Sync Definition , obwohl es kein Sync ist. Wenn mich mein nächtliches Gedächtnis nicht irrt, so ist es eine Pause.

>MS;P0=-607;P1=588;P2=380;P3=-401;P4=-9122;D=02423232323232323232323232310231010232310102323102323101023231023232323232323232323232323102323231023232323102310102310;CP=2;SP=4;R=242;O;m=2;
MS;P0=-616;P1=573;P2=378;P3=-403;P4=-9122;D=24232323232323232323232323102310102323101023231023231010232310232323232323232323232323231023232310232323231023101023102;CP=2;SP=4;R=242;
MS;P1=407;P2=-412;P4=572;P5=-598;P7=-10420;D=21712121212121212121212121245121212454512121212124542124512121212121212121212454512121212121212121212124545124512124512;CP=1;SP=7;R=72;O;
MU;P0=-401;P1=426;P3=597;P4=-582;P5=328;D=010101034101010343410101010103450103450101010501010105050343410105010101010101010103434103410103410;CP=1;R=72;

>MS;P0=-9118;P1=403;P4=-400;P6=-595;P7=599;D=1014141414141414141414141476147676141476761414761414767614147614141414141414141414141414141414141414141414141476147614;CP=1;SP=0;R=241;O;m=2;
MS;P0=-9118;P1=396;P4=-392;P6=-586;P7=601;D=10141414141414141414141414761476761414767614147614147676141476141414141414141414141414141414141414141414141414761476141;CP=1;SP=0;R=241;
MS;P0=-10016;P1=378;P2=-410;P4=574;P5=-659;D=21012121212121212121212121245121212454512121212124512124512121212121212121212454512121212121212121212124545124512124512;CP=1;SP=0;R=72;O;
MU;P0=-412;P1=399;P3=592;P4=-649;D=010101034101010343410101010103410103410101010101010101010343410101010101010101010103434103410103410;CP=1;R=73;

>MS;P0=-10454;P1=591;P2=-428;P3=401;P6=-581;P7=-320;D=3032323232323732323237323216323237161632323232321632321632323232323232323232161632373232163232321632321616161632163216;CP=3;SP=0;R=72;O;
MU;P0=388;P1=-415;P3=601;P4=-588;D=0101013401010134340101010101340101340101010101010101010134340101010134010101340101343434340134013400;CP=0;R=3;
MS;P0=-393;P1=394;P2=599;P3=-597;P4=-9116;D=314101010101010101010101010231023231010232310102310102323101023101010101010101010101010102310101023101010102310232310231;CP=1;SP=4;R=244;
MS;P0=-10500;P1=393;P3=-425;P5=-329;P6=589;P7=-634;D=1013131313131315151313151367131313676715131313156713136713131313131313131313676713131313131313131313136767136713136713;CP=1;SP=0;R=72;O;
MU;P0=415;P1=-435;P2=313;P3=581;P4=-651;P5=-333;D=010101213421010134340101010101340105340105010101010101010134340101012101010101010101343401340121340500;CP=0;R=73;

>MS;P0=-602;P2=-408;P3=386;P4=576;P5=-9116;D=23532323232323232323232323240324040323240403232403232404032324032323232323232323232323232323232323232323232323240324032;CP=3;SP=5;R=239;O;m=2;
MS;P0=-604;P2=-400;P3=392;P4=583;P5=-9116;D=35323232323232323232323232403240403232404032324032324040323240323232323232323232323232323232323232323232323232403240323;CP=3;SP=5;R=239;
MS;P1=404;P2=-411;P5=633;P6=-580;P7=-10460;D=21712121212121212121212121256121212565612121212125612125612121212121212121212565612121212121212121212125656125612125612;CP=1;SP=7;R=3;O;
MU;P0=-414;P1=401;P3=576;P4=-597;D=010101034101010343410101010103410103410101010101010101010343410101010101010101010103434103410103410;CP=1;R=73;

Bsp:
>2017.11.27 23:45:17 4: Arduino_Nano_868Mhz/msg READredu: MS;P1=407;P2=-424;P4=-327;P5=585;P6=-600;P7=-10402;D=21712121212121214121212121256121212565612121212145612125612121212121212121212565612121212121212121212125656125612125612;CP=1;SP=7;R=70;O;
2017.11.27 23:45:17 4: Arduino_Nano_868Mhz: Matched MS Protocol id 74.1 -> FS20
2017.11.27 23:45:17 3: Arduino_Nano_868Mhz: FS20 - ERROR message all bit are zeros

sidey79, 2017-11-27:
`sync => [-1,-23]`

Das ist nun halt schwer erkennbar, ob das ein Sync vor der Nachricht oder eine Pause zwischen den Nachrichten ist.
Leider sieht es halt so aus, als ob das clock Signal vor einer langen Sendepause kommt. Danach geht es ja wieder mit dem Clockpulse weiter.

HomeAutoUser, 2017-11-28:
Hallo @sidey79 
laut Beschreibung des Protokolles und auch nach den Aufzeichnungen von [diesem Signal von @elektron-bbs ](https://github.com/RFD-FHEM/RFFHEM/issues/194#issuecomment-347019566) ist es eine Pause. Ich selbst würde ein SyncPuls nur "interprätieren" wenn ich vor dem Empfang diesen Sende und mich als Empfänger dann darauf einstelle. Bei einer Wiederholung einers Signales innerhalb eines Zeitraumes "x" würde ich es als Pause definieren.

Solange wir die Erkennung nicht "stabil" hinbekommen "blähen" wir ja auch die ProtocollID´s auf mit Doppeldefinitionen, was scheinbar auch nicht unbedingt gewollt wäre. In dem Falle der Unterschiedlichen Erknnungen MC MU MS würde dies dann für ein Protocoll 3 Definitionen bedeuten.

Wie wäre es mit dem Vorschlag der Realisierung von @Ralf9 mit dem Einbau der Pause?
(Finde leider den Faden diesbezüglich nicht wieder)

sidey79, 2017-11-28:
Stimme dir zu, dass es eine Pause ist, wenn es nicht vor dem Beginn jeder Wiederholung ist.

Leider kann man das schwerlich feststellen.
Gerade wird alles ziemlich kompliziert :(

sidey79, 2017-11-28:
Irgendwie bin ich ein wenig am zweifeln, ob mein Konzept mit MU, MC und MS noch aufgeht.... sieht ja nicht so gut aus im Moment

elektron-bbs, 2017-11-28:
Mir ging schon manchmal durch den Kopf, ob es nicht besser wäre, den SIGNALduino nur die Puls-/Pausezeiten ausgeben zu lassen und die ganze Dekodierung auf dem wesentlich leistungsfähigeren Rechner durchzuführen. Das würde allerdings eine riesige Umstellung erfordern.

Einen Ansatz hätte ich erst mal noch: Die Suche nach einem Sync-Word könnte man doch sicher abbrechen, wenn es nicht gleich als erstes erscheint. Dann hat sich zumindest diese Sache schon mal erledigt.

sidey79, 2017-11-28:
Mit 2 Megabaud geht das, aber nicht mit 57600 :(

Das mit dem am Anfang kommen hat den Nachteil, dass man ja nicht weiss ob man von Beginn an mitgehört hat

elektron-bbs, 2017-11-29:
> Mit 2 Megabaud geht das, aber nicht mit 57600 :(
Man muss ja nicht unbedingt Realtime ausgeben. 2k RAM, davon 1k Array für Daten (500 int - keine Ahnung, ob noch längere Protokolle existieren), bleibt noch 1k fürs Programm.

> Das mit dem am Anfang kommen hat den Nachteil, dass man ja nicht weiss ob man von Beginn an mitgehört hat
Naja, wenn ich mir die Oszilloskope auf dieser Seite ansehe [http://tech.jolowe.se/home-automation-rf-protocol-update/](url) sollte vor dem Sync-Puls jeweils ein längerer Low-Pegel sein. Das wäre nicht der Fall, wenn der Anfang fehlt. Dann sind ja vor dem Sync normale Datenbits.

Ralf9, 2017-11-29:
> Einen Ansatz hätte ich erst mal noch: Die Suche nach einem Sync-Word könnte man doch sicher abbrechen, wenn es nicht gleich als erstes erscheint. Dann hat sich zumindest diese Sache schon mal erledigt.

Wenn bei einer MS-Nachricht der Anfang fehlt, sollte noch der erste sync der Wiederholung gefunden werden können.
Wenn man die Suche nach einem Sync-Word nach ca 100-120 Werten abbricht müsste es für MS-Nachrichten noch lang genug sein.

Gibt es bei FS20|FHT|FHTTK einen sync am Anfang?

Mich würde interessieren wie bei FS20|FHT|FHTTK die MU-Nachrichten ausschauen, wenn die MC + MS Dekoder deaktiviert sind

elektron-bbs, 2017-11-29:
Bei FS20|FHT|FHTTK gibt es weder einen Sync am Anfang, noch zwischen den Wiederholungen.
Ausführliche Tests hatte ich vor 3 Tagen, am 26.11.2017 um 13:34 gepostet.

sidey79, 2017-11-29:
@elektron-bbs 

> Naja, wenn ich mir die Oszillogramme auf dieser Seite [http://tech.jolowe.se/home-automation-rf-protocol-update/] ansehe, sollte vor dem Sync-Puls jeweils ein längerer Low-Pegel sein, auch bei Wiederholungen.

Auf der Seite des Senders trifft das zu. Auf seiten des Empfängers halt leider nicht unbedingt.
Der Empfänger dreht seine Empfindlichkeit hoch und empfängt ständig igrendwas.

Vom Prinzip ist das beschriebene Verhalten dann doch in etwa so, wie bei deaktiviertem MC und MS Dekoder. Richtig?

sidey79, 2017-11-29:
@Ralf9 

> Wenn man die Suche nach einem Sync-Word nach ca 100-120 Werten abbricht müsste es für MS-Nachrichten noch lang genug sein.

Wie meinst Du das? Wenn nach einem Sync der Darauf Folgende Sync nicht innerhalb von 100 Pulsen kommt, dann ist es ein MU Signal?

Ralf9, 2017-11-29:
Nein, wenn in den ersten ca 100-120 Pulsen kein Sync gefunden wird, dann ist es ein MU Signal

elektron-bbs, 2017-11-29:
> Der Empfänger dreht seine Empfindlichkeit hoch und empfängt ständig igrendwas.

Das sollte er eigentlich nicht. Er sollte die Empfindlichkeit knapp über den Rauschpegel einstellen und dann kommen nur ab und an Störimpulse durch. Wenn der Empfänger das nicht mehr hinbekommt, weil der Rauschteppich zu hoch ist, dann ist sowieso kein sinnvoller Empfang mehr möglich.

> Vom Prinzip ist das beschriebene Verhalten dann doch in etwa so, wie bei deaktiviertem MC und MS Dekoder. Richtig?

Diese Frage verstehe ich jetzt nicht so ganz. Welches beschriebene Verhalten meinst du?

sidey79, 2017-11-29:
Das mit dem Empfang ist aber genau so. Der Empfänger empängt ständig rauschen und der Interrupt wird ausgelöst. Daher habe ich eine Mustererkennung implementiert.
Die Sender wissen, dass der Empänger den AGC erst einstellen müssen und senden entweder eine preamble oder wiederholen das Signal einfach.

Zum beschrieben Verhalten:

> Mir ging schon manchmal durch den Kopf, ob es nicht besser wäre, den SIGNALduino nur die Puls-/Pausezeiten ausgeben zu lassen und die ganze Dekodierung auf dem wesentlich leistungsfähigeren Rechner durchzuführen. Das würde allerdings eine riesige Umstellung erfordern.

> Man muss ja nicht unbedingt in Realtime ausgeben. 2k RAM, davon 1k Array für Daten (500 int - keine Ahnung, ob noch längere Protokolle existieren), bleibt noch 1k fürs Programm.

Das ist doch in etwa die Variante des MU Decoders. Es werden die Pulszeiten ausgegeben und alles wird FHEM überlassen.

elektron-bbs, 2017-11-29:
> Das ist doch in etwa die Variante des MU Decoders. Es werden die Pulszeiten ausgegeben und alles wird FHEM überlassen.

Dann könnte man ja eigentlich MS weglassen. Der Unterschied MS/MU sollte ja eigentlich sein:
Beginn der Nachricht:
MS: Puls kurzePause Daten
MU: Daten
Zwischen Wiederholungen:
MS: Daten Pause Puls kurzePause Daten
MU: Daten Pause Daten

Zur Unterscheidung könnte dann u.U. schon die Variable "start" in der ProtocolListSIGNALduino reichen?

sidey79, 2017-11-29:
Bei MS ist der Unterschied, das die Wiederholungen nicht in Form der Signaldaten übergeben werden, sonder nur ein counter, wie viele Wiederholungen gefunden wurden.Das entlastet halt die Serielle Schnittselle... ich glaube 1 Byte dauer um die 160 uS bei 57600 Baud (aus dem Gedächtnis).  Kommt das in etwa hin....

Ralf9, 2017-11-30:
> Nein, wenn in den ersten ca 100-120 Pulsen kein Sync gefunden wird, dann ist es ein MU Signal

Ich habe es in meine Version eingebaut:
https://github.com/Ralf9/SIGNALDuino/commit/80e6be3fffb58029cc89ddbdd5c957b785cf9f12

HomeAutoUser, 2017-11-30:
Hallo, vielen Dank Ralf.
Ich habe die FW mal zum testen auf dem Gerät und parallel auch noch eine "Vorgängerversion" von dir am laufen derzeit. Ich muss auf jedenfall schonmal einen *Daumen* erteilen mit der Option des [Schwellwertes](https://github.com/Ralf9/SIGNALDuino/commit/128d57547f80a5357d971ddf2c0045ee52b5a4ac). Da ist die Auswertung auf jedenfall besser was die FS20 ... angeht. Sieht sehr gut aus. Vielleicht kann man eine Möglichkeit finden, den CSmuthresh mal automatisch zu berechnen. (vielleicht auf die ToDo und immer noch bei deaktiven MC).

Ralf9, 2017-12-01:
Das mit dem Trennen von Wiederholungen von MU-Nachrichten ist erst mal eine einfache Version zum Testen.
Das mit dem automatisch berechnen vom CSmuthresh ist wahrscheinlich nicht so einfach.
Wir können es auch mal mit einem default von 6000 versuchen oder weißt Du ein Protokoll bei dem dies nicht passen könnte?
Der einfachheit halber würde ich auch bei MU den MsMoveCountmax (MScnt) verwenden,
oder soll ich für MU einen extra MUcnt verwenden?

Es ist auch noch eine Prüfung notwendig, ob die MU-Nachricht überhaupt eine Wiederholung enthält.
Wenn die MU-Nachricht keine Wiederholung enthält, dann muß sie auch nicht nach der Pause durchsucht werden.
Einige MU-Nachrichten sind recht lang wie z.B. das Protocol 9.

Eine Möglichkeit ist das pattern Array nach Werten größer muthresh zu durchsuchen. 
Wenn der gefundene Wert einmal vorkommt, dann wird geschaut ob dieser Wert in der ersten 10 Pulsen vorkommt. Wenn ja, dann kann es kein Wiederholungstrenner sein.
Wenn der gefundene Wert zu oft (ca 5-10 mal?) vorkommt, dann ist es höchtwahrscheinlich auch kein Wiederholungstrenner.

elektron-bbs, 2017-12-03:
Ich habe mich jetzt mal mit dem hier beschriebenen Fehler näher auseinander gesetzt:
[https://github.com/RFD-FHEM/SIGNALDuino/issues/76#issuecomment-347007613](url)

Ich habe sämtliche commits im genannten Zeitraum durchprobiert. Der genannte Fehler tritt auf mit dem 2. commit am Nov 20, 2017 (Raised SignalDetectorClass version to 1.1.0 and added ms sync detect fix) [https://github.com/RFD-FHEM/SIGNALDuino/commit/8ca3123ee80ac0c7b21930e22078efaa3bb655c7](url) und zieht sich durch bis zur aktuellen Version.

Wenn ich in der aktuellen Version diese Änderung rückgängig mache, wird der FHT-Temperaturregler wieder dekodiert:

 @@ -985,7 +985,7 @@ bool SignalDetectorClass::getSync() 
985 985    				(syncabs > syncMinFact*pattern[clock]) &&  
986 986    				// (syncabs < maxPulse) &&  
987 987    				//	 (validSequence(&pattern[clock],&pattern[p])) &&  
988    -				(histo[p] < messageLen*0.08) && (histo[p] > 1)  
 988   +				(histo[p] < messageLen*0.08) && (histo[p] >= 1)  
989 989    				//(histo[p] < 8) && (histo[p] > 1)  
990 990      

Anbei noch meine Notizen und Logs dazu: 
[Bug_2017-12-02.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/1524716/Bug_2017-12-02.txt)

HomeAutoUser, 2017-12-28:
Hallo,
ich möchte mal noch hieran erinnern ;-)

>2017.12.28 23:13:23 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-388;P1=400;P3=610;P4=-620;P5=-8972;D=0101010101010101010101010341010103410343410341010103410343434101010101010101010103410341010343410101010101034343410343434341034341034101510101010101010101010101034101010341034341034101010341034343410101010101010101010341034101034341010101010103434341034;CP=1;R=15;O;
2017.12.28 23:13:23 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.28 23:13:28 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0

>2017.12.28 23:14:14 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=393;P2=-400;P3=610;P4=-614;P5=-8964;D=0121212121212121212121212341212123412343434121234121212121212341212121212121212123412341212343412121212121212121212121212121234121234121512121212121212121212121234121212341234343412123412121212121234121212121212121212341234121234341212121212121212121212;CP=1;R=254;O;
2017.12.28 23:14:14 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate

Keine Trennung, ein Overflow und somit auch keine Verarbeitung weil dies außerhalb der Längenprüfung ist. @sidey79 was sprach gegen die Lösung von Ralf mit seiner Trennung dies zu übernehmen, "außer" das man für den "festen Wert" eine Automatik schaffen müsste.

HomeAutoUser, 2017-12-28:
@sidey79,
mir ist auch aufgefallen soeben dies.

>2017.12.28 23:21:29 4: Arduino_Nano_868Mhz/keepalive ok, retry = 0
2017.12.28 23:21:43 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=401;P2=-387;P3=606;P4=-599;P5=-8960;D=0121212121212121212121212341212341212121212341212341234341212341212121212121212121212341212343412341212121234343434123412121234341234121512121212121212121212121234121234121212121234121234123434121234121212121212121212121234121234341234121212123434343412;CP=1;R=40;O;
2017.12.28 23:21:43 4: Arduino_Nano_868Mhz: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.28 23:21:43 5: Arduino_Nano_868Mhz: Starting demodulation at Position 1
2017.12.28 23:21:43 5: Arduino_Nano_868Mhz: applying postDemodulation
2017.12.28 23:21:43 5: Arduino_Nano_868Mhz: modified value after postDemodulation:1
2017.12.28 23:21:54 4: Arduino_Nano_868Mhz/msg READredu: MU;P0=-32001;P1=392;P2=-396;P3=599;P4=-620;P5=-8960;

Diese Meldung würd mit dem "1." Protokoll verglichen und dann gehts nicht weiter, OBWOHL es eine gültige Definition wäre für das Protokoll 73 FHT. Hast du dort ein "HALT" verbaut weil dies ein Overflow ist?

sidey79, 2017-12-28:
@HomeAutoUser 

Leider gibt es von Ralfs Anpassung wenig Dokumentation. Er hat mittlerweile so viele Änderungen vorgenommen, dass es nicht mehr einfach möglich ist, den Hintergrund einer Änderung zu verstehen um entscheiden zu können, ob ich diese übernehmen soll.
https://github.com/RFD-FHEM/SIGNALDuino/compare/dev-r33-cc1101-mcfirtbitfix...Ralf9:dev-r332_cc1101

Was das Trennen angeht, bin ich noch unsicher ob das der richtige Weg ist. Wenn ich seinen Ansatz korrekt verstanden habe, dann verwirft er einfach alles vor einer Pause und hofft dass dann eine 2. Pause gleicher Länge noch in den Puffer passt :)

Vom Prinzip ähnelt das dem Vorgehen bei dem MS Signalen , wobei wir da eine Kombination Puls + Pause suchen.
Das Prinzip eines MU Signales ist aber eben, dass es keinen Sync oder Start gibt den wir kennen.

Ich habe dazu bereits zwei überlegungen angestellt:
1.  https://github.com/RFD-FHEM/SIGNALduino/compare/dev-r33-cc1101-mcfirtbitfix...experimental-mu
 
-> Hier wird die Ausgabe fortgesetzt und nicht beendet. Klappt gut, solange die Datenrate nicht höher als die Ausgabezeit ist. Klappt prinzipiell, aber leider nicht immer.

2. https://github.com/RFD-FHEM/SIGNALduino/compare/dev-r33-cc1101-mcfirtbitfix...experimental-syncprotocols

-> Wenn zweimal ein langer pausepuls im Puffer ist, wird es als MS Signal gewertet. Trifft leider in der Regel nur zu, wenn die Signalfolge kurz genug ist um in den Puffer zu passen. 

Was deine Feststellung mit Protokoll 73 angeht tippe ich, dass Du die ID 73 nicht im Attribut development angegeben hast.
`attr development "73y"`

Abgesehen von diesem Thema, bin ich dabei die SIGNALdecoder Klasse für ESP und Arduino zu vereinheitlichen.

elektron-bbs, 2017-12-29:
> -> Wenn zweimal ein langer pausepuls im Puffer ist, wird es als MS Signal gewertet. Trifft leider in der Regel nur zu, wenn die Signalfolge kurz genug ist um in den Puffer zu passen.

Das halte ich nicht für einen brauchbaren Ansatz. Das würde zwar bei FS20 passen (Nachricht-Pause-Nachricht-Pause-Nachricht), aber nicht bei FHT (Nachricht-Pause-Nachricht). Beide Protokolle sind identisch, bis auf die Anzahl der Wiederholungen. Dann müsste aber eines als MS und eines als MU definiert werden.

Worin besteht eigentlich das Problem, die Nachrichten nach einem Low-Pegel, der länger ist als z.B. 10*longlow zu trennen? Nach einer Pause länger als 32 mSek geht es doch auch.

sidey79, 2017-12-29:
Danke für den Hinweis.

10*longlow könnte man nur bei Manchester anwenden. Sonst kennen wir kein longlow.
Außerdem gibt es da einen Rauchmelder, der hatte eine Pause um die 15ms. Der müsste berücksichtigt werden.

elektron-bbs, 2017-12-30:
"longlow" war wohl etwas missverständlich ausgedrückt. Der SIGNALduino sammelt doch Puls-/Pausenzeiten. Diese betragen z.B. 400/-400 für eine 0 und 600/-600 für eine 1 bei FS20/FHT und FHTTK. Der längere Low-Pegel wäre hier z.B. -600. Diesen mal 10 ergibt -6000, ab dem eine Trennung erfolgen könnte. Die Pausenzeit zwischen den Wiederholungen ist normalerweise -10000.

sidey79, 2017-12-30:
Ich verstehe. Man könnte versuchen das über einen Durchschnitt zu ermitteln. Weicht der neue Wert mehrfach vom Durchschnitt ab, dann soll ProcessMessage ausgeführt werden.

elektron-bbs, 2017-12-30:
Die Durchschnitte der Puls-/Pausezeiten müssten ja eigentlich schon ermittelt werden. Oder was wird am Ende bei P0, P1,P2 etc. ausgegeben?

sidey79, 2017-12-30:
Der Durchschnitt aller Pulse wird noch nicht errechnet.
Könnte man ja als moving average implementieren. Damit würden dann halt aber Wiederholungen nicht mehr als Wiederholung erkannt.

elektron-bbs, 2017-12-30:
Na, doch, sobald die Pause zehnmal größer als der Durchschnitt  ist, wird getrennt und dann von vorn angefangen.

sidey79, 2017-12-30:
Genau, bei MS Protokollen wird die Anzahl an Wiederholungen mitgezählt aber nicht jede Wiederholung separat übergeben.  Dieses Verhalten wäre dann weg. Eventuell könnte man hier was mit einer einfachen Checksumme machen. Das hatte ich vor langer Zeit schon Mal überlegt dann aber verworfen.

elektron-bbs, 2017-12-30:
Bei FS20/FHT/FHTTK müsste nicht jede Wiederholung ausgegeben werden, da sie gleich sind. Ich glaube mich aber zu erinnern, das bei Siro die 2. Nachricht wichtig ist, da sie andere Daten enthält.

sidey79, 2018-02-18:
Ich versuche das Fehlerfahalten, "Falsche MC Erkennung" nachzustellen.

Allerdings habe ich bis jetzt noch kein Signal gefunden, mit dem ich den Fehler nachstellen konnte. :(
Die DMC Ausgabe war ja auch lange zeit fehlerhaft.

Hat jemand ein MU Signal für mich, das bei aktiviertem MC Decoder als MC erkannt wird?

HomeAutoUser, 2018-02-18:
>Hat jemand ein MU Signal für mich, das bei aktiviertem MC Decoder als MC erkannt wird?

Dann werde ich mal den MC aktivieren und nachsehen.

HomeAutoUser, 2018-02-18:
>Du könntest die DMC Ausgabe noch mal in der FW aktivieren...

Da muss ich nochmal anfragen wie ich diese aktiviere. *liebschau*

sidey79, 2018-02-18:
Schau nach dem Abschnitt

```#if DEBUGDECODE == 1 // todo kommentar entfernen

					SDC_WRITE(MSG_START);

					SDC_PRINT("DMc");

~```

HomeAutoUser, 2018-02-18:
Habe es gefunden.

>2018.02.18 19:16:19 4: nano_868Mhz/msg READ: DMc;P0=418;P1=-1044;P2=-567;P3=908;D=01023202020102020202320202020131020232020241020231313131023132020201025;
2018.02.18 19:16:19 4: nano_868Mhz/msg READ: Mc;LL=-1044;LH=908;SL=-567;SH=418;D=61F05C0;C=489;L=25;R=213;

>2018.02.18 19:17:33 4: nano_868Mhz/msg READ: DMc;P1=592;P2=-412;P3=407;P6=-592;D=12323232323232323232323216323232161632323232321632321632323232323232323232323232323232163232323216321616163216163216303232323232323232323232321632323216163232323232163232163232323232323232323232323232323216323232321632161616321616321630323232323232323232;O;
2018.02.18 19:17:33 4: nano_868Mhz/msg READ: Mc;LL=-592;LH=592;SL=-412;SH=407;D=FFF7AFDDFFFF7DAB5A;C=333;L=71;R=74;

Hier sind 3 MC und wenn ich das richtig sehe, sind es die 3 Teilungen der MU Nachricht welche als MC defineirt werden.
>2018.02.18 19:17:49 4: nano_868Mhz/msg READ: DMc;P1=922;P2=-1047;P3=-537;P4=434;D=0121213434243434312434313424312134342124312124313424313434342434343431343434342124343134343424343121212124312134343434243121342134243;
2018.02.18 19:17:49 4: nano_868Mhz/msg READ: Mc;LL=-1047;LH=922;SL=-537;SH=434;D=A8F7345ACC3E0B8755A0D26;C=489;L=91;R=216;
2018.02.18 19:17:49 4: nano_868Mhz/msg READ: DMc;P1=909;P2=-1043;P3=-543;P4=430;D=0121213434243434312434313424312134342124312124343124313434342434343431343434342124343134343424343121212124312134343424343134342121243;
2018.02.18 19:17:49 4: nano_868Mhz/msg READ: Mc;LL=-1043;LH=909;SL=-543;SH=430;D=A8F7345AEC3E0B8755A1C56;C=487;L=91;R=217;
2018.02.18 19:17:49 4: nano_868Mhz/msg READ: DMc;P1=923;P2=-1018;P3=-536;P4=451;D=0121213434243434312434313424312134342124312124312431343434243434343134343434212434313434342434312121212431213434343434212434343431243;
2018.02.18 19:17:49 4: nano_868Mhz/msg READ: Mc;LL=-1018;LH=923;SL=-536;SH=451;D=A8F7345AD87C170EAB40BEC;C=487;L=90;R=216;

sidey79, 2018-02-18:
Top,

`		std::string dstr = "DMc;P1=922;P2=-1047;P3=-537;P4=434;D=121213434243434312434313424312134342124312124313424313434342434343431343434342124343134343424343121212124312134343434243121342134243;";
`

Damit klappt es den Fehler nachzustellen :)

elektron-bbs, 2018-02-18:
Hat nur einen Haken: Diese Nachrichten sind invertierte Hideki-Übertragungen, erkennbar an den Puls-/Pausezeiten von ca. 1000 und 500 und dem Beginn mit hex A8.
Die FS20, FHT und FHTTK haben Zeiten von ca. 600 und 400 und beginnen mit einer Präambel mit 12 Nullen. Wenn diese als MC verarbeitet werden beginnen diese meistens mit hex FFF, wie in diesem Beispiel von @HomeAutoUser

HomeAutoUser, 2018-02-18:
@sidey79 
>2018.02.18 21:07:52 4: nano_868Mhz/msg READ: DMc;P0=-394;P1=397;P4=-603;P6=593;D=12343432546464101064641064106464106464641010646410101010101010101010106410646410646410106410101;
2018.02.18 21:07:52 4: nano_868Mhz/msg READ: Mc;LL=-603;LH=593;SL=-394;SH=397;D=51494A8A00129440;C=331;L=61;R=242;

hier ist eine FS20 Zeit wo MC herasukommt.

[41 more replies]

commit 93243a40 (2018-02-24): signalDecoder.cpp

In ManchesterpatternDecoder isManchester wird geprüft, ob longlow + longhigh und longhigh+longlow als Sequenz vorkommen.
Dies ist notwendig, um eine bessere MC Erkennung zu erreichen. In der Regel, sollten MC Signale auch beide Sequenzen enthalten. Auch wenn dies nicht 100% garantiert werden kann. z.B bei FFFF ist dies nicht der Fall. Dieser Fall wird aber auch heute bereits nicht erkannt.

fixes #76, #72
closed 2019-12-03 by sidey79