RFD-FHEM/SIGNALDuino #78: CUL_WS - WS7000 (433Mhz) Erkennung ab Commit 171115 fehlerhaft

HomeAutoUser, 2017-11-25:
Hallo @sidey79,
seit Tagen hatte ich keinen Empfang des WS7000. Heute habe ich mich mal auf die Schliche gemacht woran es liegt. Meine erste Vermutung, die Batterien waren es nicht. Dann habe ich sämtlicht FW Versionen von der aktuellsten bis hin zum [Commits on Nov 14, 2017 ](https://github.com/RFD-FHEM/SIGNALDuino/tree/a03b1d6aefe0b81604023a3153759f3084e23c35) down gegradet. Dort ist die Erkennung sofort wieder da und im normalen Rhytmus. Bitte umgehend prüfen!

RAWMSG - mit FW V 3.3.1-dev_171114 SIGNALduino lt. o.g. Link
> MU;P0=-937;P1=774;P2=-448;P3=283;D=012121212121212121212301212301230303030123030301212301212301230123012123030121230301212301230121230123012121212303012121230123030123012303030301230301230301230123;CP=3;R=38;

elektron-bbs, 2017-11-25:
Die Puls-/Pausezeiten weichen schon ziemlich stark ab von den geforderten:
	    # kurz 366 µSek / lang 854 µSek / gesamt 1220 µSek - Sollzeiten

sidey79, 2017-11-25:
Hallo @HomeAutoUser 

Auf Anhieb kann ich nichts feststellen, was diesen Fehler verursacht haben könnte.
Einzig 6414e4fbd60901ee51e61e3c0fefb7ed2be7bae6 könnte sich ausgewirkt haben, alles andere war Manchester Bezogen.

Gibt es denn irgendwelche Ausgaben, welche Du dem Sensor zuordnen könntest?

Ralf9, 2017-11-25:
@HomeAutoUser
funktioniert es, wenn Du den Manchester Decoder deaktivierst?

HomeAutoUser, 2017-11-26:
Hallo,
ich habe nun nochmal getestet.

Vorgehen:
Sobald ich **MC deaktiviere**, funktioniert dieser sofort wieder.
Batterrien rein raus.... mehrere Sendungen und Werte.
- - - -- - - -- - - -- - - -- - - -- - - -
MC aktiviert
Batterrien rein raus.....
kein Empfang
Batterrien rein raus.....
kein Empfang
- - - -- - - -- - - -- - - -- - - -- - - -
MC deaktiviert, sofort erhalte ich Nachrichten

**Bei aktiven MC wird der Sensor neuerdings mal als MS erkannt und mal als MU.** *Augen roll*

>2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ: MS;P2=-3884;P3=508;P4=-1919;P5=-951;D=32343535353535353535353535353535353535343434343535343434343535353534353534;CP=3;SP=2;R=220;O;m2;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ: MS;P2=-3886;P3=520;P4=-1923;P5=-948;D=32343535353535353535353535353535353535343434343535343434343535353534353534;CP=3;SP=2;R=220;O;m1;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ: MS;P2=-3886;P3=510;P4=-1929;P5=-941;D=32343535353535353535353535353535353535343434343535343434343535353534353534;CP=3;SP=2;R=220;m0;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-1228;P1=-232;P2=744;P3=510;P4=-1929;P5=-941;P6=996;P7=-476;D=343535353535353535353535353535673535343430313031352;CP=3;R=220;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-712;P1=754;P2=-1827;P3=-938;P4=514;P5=-214;P6=-1210;D=012121345454645454310101345454545454543104245454645;CP=4;R=224;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=10  seq_even[2]=23  seq_odd[2]=30  seq_even[3]=103  seq_odd[3]=34MU;P2=-464;P7=756;D=01212121212101230343510123032151510121062727272727272727272;CP=1;R=61;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=32  seq_even[2]=-1  seq_odd[2]=30MU;P0=281;P1=-941;P2=772;P3=-447;D=012323012301010101230101012323012301232301230123230123230123010123230101010123230101012323010123232301230101230123010101012323012301010123230;CP=0;R=54;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=32  seq_even[2]=23  seq_odd[2]=30  seq_even[3]=104  seq_odd[3]=40MU;P0=509;P2=986;P3=-503;P4=-936;D=012123230404030404040304030404230404040404040323242323230404032423040505040324232323040403050304032423230403040404030405230404040404040404032324232323040403242121212423232304040324250404032424242304037;CP=0;R=227;
2017.11.26 13:21:53 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-292;P1=503;P2=-965;P3=-1941;D=012131312121212121212121312131212131313131313121212;CP=1;R=224;
2017.11.26 13:21:54 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-1929;P1=504;P2=-965;P3=-4001;P4=390;D=13401012101012121010121212121212121210121012121010101010101212121212121212;CP=1;SP=3;R=223;O;m2;
2017.11.26 13:21:54 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-1937;P1=496;P2=-952;P3=-3954;D=13101012101012121010121212121212121210121012121010101010101212121212121212;CP=1;SP=3;R=223;m1;
2017.11.26 13:21:54 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-1937;P1=496;P2=-952;P3=-3954;D=13101012101012121010121212121212121210121012121010101010101212121212121212;CP=1;SP=3;R=223;m0;
2017.11.26 13:21:54 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-1937;P1=496;P2=-952;P3=-3954;D=13101012101012121010121212121212121210121012121010101010101212121212121212;CP=1;SP=3;R=223;
2017.11.26 13:21:56 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1MU;P1=766;P2=-461;P3=279;P4=-949;D=012121212121212121212341212341234343434123412123412341234121234123412123412123412343412123434343412123434121212341212121234121212343412343434341234343434341234343;CP=3;R=54;
2017.11.26 13:21:57 4: Arduino_Nano_433Mhz/msg READ: MS;P1=475;P2=-7395;P3=-2029;P4=-3934;D=12131413131414141413131313131413131413131313141413;CP=1;SP=2;R=237;O;m2;
2017.11.26 13:21:58 4: Arduino_Nano_433Mhz/msg READ: MS;P1=459;P2=-7395;P3=-2029;P4=-3934;D=12131413131414141413131313131413131413131313141413;CP=1;SP=2;R=237;O;m1;
2017.11.26 13:21:58 4: Arduino_Nano_433Mhz/msg READ: MS;P1=457;P2=-7395;P3=-2029;P4=-3934;D=12131413131414141413131313131413131413131313141413;CP=1;SP=2;R=237;O;m0;
2017.11.26 13:21:58 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-1981;P1=468;P2=-3950;P4=-7958;D=14101210101212121210101010101210101210101010121210;CP=1;SP=4;R=237;O;m2;
2017.11.26 13:21:59 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-2028;P1=471;P2=-3935;P4=-7963;D=14101210101212121210101010101210101210101010121210;CP=1;SP=4;R=237;O;m1;
2017.11.26 13:21:59 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-2020;P1=454;P2=-3943;P4=-7969;D=14101210101212121210101010101210101210101010121210;CP=1;SP=4;R=237;O;m0;
2017.11.26 13:21:59 4: Arduino_Nano_433Mhz/msg READ: MS;P0=-2017;P1=461;P2=-3940;P4=-7972;D=14101210101212121210101010101210101210101010121210;CP=1;SP=4;R=237;O;
2017.11.26 13:21:59 4: Arduino_Nano_433Mhz/msg READ: MS;P0=456;P1=-3952;P2=-2002;P3=-7964;D=03020102020101010102020202020102020102020202010102;CP=0;SP=3;R=238;O;m2;
2017.11.26 13:21:59 4: Arduino_Nano_433Mhz/msg READ: MU;P0=456;P4=-931;P5=771;P6=-452;D=030201020201010456565656565656565656745656745674747474567456567456745674565674567456567456567456747456567474747456567474565656745656565674565656747456747474745674747474745674747;CP=7;R=238;
2017.11.26 13:21:59 4: Arduino_Nano_433Mhz: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
[... cut ...] 4343434123434121212343434121234123412123434341212341212123434343412123434121212341212121234121212343412341234341212123434343434343;CP=3;R=57;
2017.11.26 13:22:48 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=21  seq_odd[1]=34  seq_even[2]=23  seq_odd[2]=-1MU;P1=765;P2=-455;P3=277;P4=-941;D=012121212121212121212341212341234343434123412341212343434121234123412123434341212341212123434343412123434121212341212121234121212343434341234341234123434343412123;CP=3;R=57;

HomeAutoUser, 2017-11-26:
@sidey79 

Ich hoffe nun die richtigen Nachrichten gefiltert zu haben. Das Problem ist bei mir, ich kann das Zeitlich schwer eingrenzen bzw filtern weilbei mir "ständig mindestens ein Sensor sendet". Da würde ich mir manchmal einen "anderen Filter" wünschen um nur den einen Sensor zu erkennen. [Anderes Thema aber]

Ich habe nun die gefixte FW genutzt und bei aktiven MC wird der Sensor nicht erkannt.

>2017.11.26 16:56:06 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-945;P1=766;P2=-447;P3=281;D=012121212121212121212301212301230303030123012303012303012121230123012123012121230301230121230121230123012301212301212121230121212303012303030301212303030303012303;CP=3;R=40;
2017.11.26 16:56:16 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-2276;P1=774;P2=-451;P3=281;P4=-931;D=012121212121212121212341212341234343434123412121234341234121234341234123412121212343412341234341212343412121234343434121234123412123412121234343412121234343412123;CP=3;R=39;
2017.11.26 16:59:02 4: Arduino_Nano_433Mhz/msg READ: MU;P0=-32001;P1=779;P2=-460;P3=276;P4=-951;D=012121212121212121212341212341234343434123434343412343412121234123412123412121212343412341234341212343412343412341234121234121234123434121212341234123434343434343;CP=3;R=37;

>2017.11.26 17:00:34 4: set Arduino_Nano_433Mhz enableMessagetype manchesterMC CEC

>2017.11.26 17:09:37 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=32  seq_even[2]=21  seq_odd[2]=30  seq_even[3]=103  seq_odd[3]=10DMc;P0=456;P1=-522;P2=935;P3=-1019;D=0123232301230123210103230123230101232101030101012
2017.11.26 17:09:37 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-1019;LH=935;SL=-522;SH=456;D=54974A2E1;C=488;L=36;R=37;
2017.11.26 17:09:37 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=12  seq_odd[0]=23  seq_even[1]=34  seq_odd[1]=45  seq_even[2]=54  seq_odd[2]=46  seq_even[3]=67  seq_odd[3]=71DMc;P0=-529;P1=446;P3=939;P7=-1012;D=12345467103737371037103730101737103737103710301017101010373010101010173730101017103017371010373710301710101030171030101010171010
2017.11.26 17:09:37 4: Arduino_Nano_433Mhz/msg READ: Mc;LL=-1012;LH=939;SL=-529;SH=446;D=2A4BA5270BF5E68A619F0;C=487;L=83;R=3;
2017.11.26 17:09:37 4: Arduino_Nano_433Mhz/msg READ:   seq_even[0]=101  seq_odd[0]=12  seq_even[1]=23  seq_odd[1]=30  seq_even[2]=21  seq_odd[2]=32  seq_even[3]=103  seq_odd[3]=-1MU;P0=-925;P1=779;P2=-448;P3=279;D=0123012123212121212301212301230303030123012121230301230121230303030121230123012123012121230303012121230303012123;CP=3;R=42;

Ralf9, 2017-11-26:
> Das Mit der MS Erkennung könnte eigentlich nur an dieser Anpassung liegen:
> 8ca3123
> 
> Meist haben die MS Nachrichten aber Wiederholungen und wären somit ohne die Änderung auch als MS erkannt worden.

Ja dies ist mir auch schon aufgefallen, dies betrifft aber nur MS Nachrichten ohne Wiederholungen.
Ist es denkbar, daß es sowas geben könnte?
Diese wurden seither als MU-Nachrichten erkannt.

Z.B. bei Nachrichten mit der Protokoll ID 7 wurden seither die letzte Wiederholung als MU ausgegeben:
```

MS;P0=-3892;P1=492;P2=-971;P3=-1958;D=10121313121312131313121312121212121313121212121312131313131212131312131313;CP=1;SP=0;R=0;O;m13;
MS;P0=-3889;P1=490;P2=-980;P3=-1936;D=10121313121312131313121312121212121313121212121312131313131212131312131313;CP=1;SP=0;R=0;O;m12;
MS;P0=-3888;P1=489;P2=-972;P3=-1960;D=10121313121312131313121312121212121313121212121312131313131212131312131313;CP=1;SP=0;R=0;O;m11;
MS;P0=-3892;P1=479;P2=-984;P3=-1942;D=10121313121312131313121312121212121313121212121312131313131212131312131313;CP=1;SP=0;R=0;O;m10;
MS;P0=-3892;P1=485;P2=-985;P3=-1957;D=10121313121312131313121312121212121313121212121312131313131212131312131313;CP=1;SP=0;R=0;m9;
MS;P0=-3892;P1=485;P2=-985;P3=-1957;D=10121313121312131313121312121212121313121212121312131313131212131312131313;CP=1;SP=0;R=0;m8;
MU;P0=-3892;P1=485;P2=-985;P3=-1957;P7=240;D=101213131213121313131213121212121213131212121213121313131312121313121313137;CP=1;R=0;e;
```

sidey79, 2017-11-26:
@HomeAutoUser

Da sind immer noch die Debug Ausgaben enthalten.
Hast Du debugdetect definiert?

HomeAutoUser, 2017-11-26:
Das einzige was bei mir definiert ist in den Attributen ist Debug=0 und verbose=4. Mir ist nicht bewusst weiter irgendwo debugdetect definiert oder abgeändert zu haben.

sidey79, 2017-11-26:
@HomeAutoUser 
Das ist seltsam, da seit https://github.com/RFD-FHEM/SIGNALDuino/commit/d0fa27131005043c7b8e7d604407e90e6a3cdef6 die Ausgabe nur noch erfolgt  wenn debugdetect definiert wurde.

Erkennbar an so einer Ausgabe:
`
2017.11.26 17:09:37 4: Arduino_Nano_433Mhz/msg READ: seq_even[0]=101 seq_odd[0]=12 seq_even[1]=23 seq_odd[1]=30 seq_even[2]=21 seq_odd[2]=32 seq_even[3]=103 seq_odd[3]=-1�MU;P0=-925;P1=779;P2=-448;P3=279;D=0123012123212121212301212301230303030123012121230301230121230303030121230123012123012121230303012121230303012123;CP=3;R=42;�`

HomeAutoUser, 2017-11-27:
@sidey79 
Ich habe nun den aktuellen [Commit](https://github.com/RFD-FHEM/SIGNALDuino/tree/736164fb9de6ecbd89d4ad2b75be89b6d73a67ef) ausprobiert.
Der Empfang des WS7000 ist wieder vorhanden. Die Ausgaben
>READ: seq_even[0]=101 seq_odd[0]=12 seq_even[1]=23 seq_odd[1]=30 seq_even[2]=21 seq_odd ..

sehe ich auch nicht mehr. Ich behalte dies im Auge.

sidey79, 2017-11-27:
@Ralf9

Naja unwahrscheinlich, allerdings habe ich beim Testen festgestellt, dass nicht immer zwei Wiederholungen im Speicher sind. Das passiert sogar bei itv3 .

Durch die Anpassungen wird das deutlich verbessert.

Denkbar wäre natürlich, dass jemand auf Wiederholungen verzichtet und vor der Nachricht eine andere Aufwecken Sequenz sendet.

HomeAutoUser, 2018-04-05:
Ich beende dies Thema mal, da der CUL_WS WS7000 ja empfangen wird bzw. bei aktuellen Problem in dem dazu erstellen Thema geschrieben wird.

closed 2018-04-05 by HomeAutoUser