RFD-FHEM/SIGNALDuino #79: Lange Protokolle / MU Ausgabe

sidey79, 2017-12-02:
Hier wird ein Protokoll beschrieben, welches aus ca. 244 Pulsen besteht:

https://forum.fhem.de/index.php?topic=80164.0

Es handelt sich um ein MU Protokoll, welches unter Umständen als MC erkannt werden könnte.
Das ist erst mal Nebensache.

Die Problematik besteht eher darin, dass es unwahrscheinlich ist, dass eine Übertragung komplett im Puffer liegt.

Es gäbe zwei Optionen:
1. Puffer vergrößern, in diesem Fall müsste er auf das doppelte der aktuellen Größe anwachsen. 
2. Bei MU Protokollen, die aufgrund von vollem Puffer ausgegeben werden könnte man auch überlegen die serielle Übertragung nicht zu beenden und wenn mehr Daten vorliegen diese zu übermitteln.

sidey79, 2017-12-02:
@ralf9 was meinst Du?

Ralf9, 2017-12-02:
Ich sehe da kein Problem. Eine Nachricht besteht aus ca 225 Pulsen und danach kommt die Wiederholung. 
Durch den aktivierten debug ist es sehr unübersichtlich. Mit verbose 4 wäre es deutlich übersichtlicher.

sidey79, 2017-12-02:
Der Empfänger bekommt die Übertragung nicht von Beginn an mit. Somit sind nicht zwei Startsequenzen im Puffer.

Man hat das Ende von einer Übertragung und den Beginn einer weiteren.

Ralf9, 2017-12-02:
Bei dieser Nachricht ist der Begin im Puffer

```
MU;P0=156;P1=-1764;P2=290;P3=-798;P4=657;P5=-432;P6=-11396;P7=5224;D=01234523232345452323452323232323234545234523454545232323232323234523452345234523452323234523232345232323232345452323232323232323232323234523454523232323454545452323232323452345232323452345234545232345454523234523454523454523267123452323234545232345232323;CP=2;R=107;O;
``` 

```
 01 2345232323454523234523232323232345452345234545452323232323232345234523452345234523232345232323452323232323454523232323232323232323232345234545232323234545454523232323234523452323234523452345452323454545232345234545234545232
671 23452323234545232345232323;
```

sidey79, 2017-12-02:
Stimmt, aber die startsequenz fehlt am Anfang. Würde man danach suchen, findet man das signal nicht

sidey79, 2017-12-03:
Seltsam, ich dachte ich hätte heute hier etwas gepostet.... 

Ich habe ein wenig experimentiert:

Ich habe ein Flat: muDetected nach der gleichen Methode die mcDetected implementiert.
Es wird nur gesetzt, wenn ein overflow (Puffer voll) erkannt wurde.

Die Ausgabe der Pulse P1=x;P2=y; habe ich an das Ende der seriellen Übertragung geschoben.
Die Ausgabe beginnt dann mit MU;D=121212. Somit kann die Ausgabe (D) fortgesetzt werden.
Geht prinzipiell auch, aber irgendwie werden die Pulse (P) unter umständen aus dem pattern array entfernt und sind dann nicht mehr alle enthalten.
Genauer untersucht habe ich den Fall noch nicht, wieso Pulse überschrieben werden. Man würde das Problem vermutlich in den Griff bekommen.

Ein anderer Ansatz wäre,  wie bisher MU;P1=x;P2=y;D=121212 zu übergeben, wird overflow (puffer voll) erkannt, dann wird der Nächste Pufferinhalt auch ausgegeben. Das MU können wir uns schenken, die Pulse und die Signaldaten  P1=x;P2=y;D=121212; brauchen wird.

Das sähe dann in etwa so aus:
<STX>MU;P1=x;P2=y;D=121212;P1=x;P2=y;D=121212;cp=1;o;<ETX>

Diese Daten könnte man in FHEM dann zusammensetzen. Geht mit ein bisschen Fleßarbeit.
Diese Variante hätte den Vorteil, dass es auch mit der message reduction funktioniert. Da ist ja unklar, ob wir eine gerade Anzahl an Daten oder ungerade haben. :(

Eine dritte alternative wäre, dass unseren Puffer nicht komplett leeren.
Nehmen wir an, wir haben die Daten mit einem Startpuls (>5000 uS) im Puffer.
Wir geben den kompletten Puffer an FHEM aus. Sollten wir machen, da wir nicht wissen was wir wirklih haben. Heute löschen wir den Puffer danach komplett. Wir könnten auch ein buffermove bis zu dem ersten gefunden Startpuls (>5000) machen. Einen Teil der Daten geben wir dann doppelt aus, so müssten aber Protokolle bis 253 Datenpulse in den Puffer passen.

@Ralf9 
Variante 3 müsste sich doch mit dem mucut irgendwie vereinen lassen oder?

Ralf9, 2017-12-03:
`MU;P0=156;P1=-1764;P2=290;P3=-798;P4=657;P5=-432;P6=-11396;P7=5224;D=01234523232345452323452323232323234545234523454545232323232323234523452345234523452323234523232345232323232345452323232323232323232323234523454523232323454545452323232323452345232323452345234545232345454523234523454523454523267123452323234545232345232323;CP=2;R=107;O;`

Mit dem Mucut müsste es eigentlich passen.
Die Nachricht wird bis P6 ausgegeben
`012345232323454523234523232323232345452345234545452323232323232345234523452345234523232345232323452323232323454523232323232323232323232345234545232323234545454523232323234523452323234523452345452323454545232345234545234545232`

Dann wird ein buffermove bis P6 gemacht
`67123452323234545232345232323`
und dann weiter empfangen.

Ich habe den Mucut und die Ausgabe der Wiederholungen inzwischen in meiner Version eingebaut.

sidey79, 2017-12-03:
Muss noch ergänzen. Variante drei hilft mir nicht bei meinem Lichterkerzen Protokoll.
Da ist der Trenner zwischen den Wiederholungen knappe 100uS ...

Habe mal eben Variante 2 ausprobiert (ohne reduction)

`MU;;P0=-182;P1=129;P2=434;P3=98;D=01020202020101010101010101010101010101010101010101010101010101010101010101010103010101010101010101010101010101010101010101010101010101010102020202010101010101010101010101010101010101010101010101010101010101010101010101010103010101010101010101010101010101;P0=-178;P1=130;P2=439;P3=98;P5=164;D=01010101010101010101020202020101010101010101010101010101010102020202010501010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010202020201010101010101010101010101010101010101010101010101;P0=-179;P1=126;P2=441;P3=98;P5=164;P6=-152;D=01010101010101010101010101010101010101010101010101020201010101020202020101010102020202010101010202020201010101020202020101010102020202010101010202020201010101020202020101010102620202010101010202020201010101020202020101010102020202010101010202020201010101;P0=-188;P1=120;P2=433;P3=98;P5=164;P6=-152;D=02020202010101010202020201010101;CP=1;R=27;o;`

Vemutlich fehlen hier aber Daten, da der FIFO schneller volläuft, als die ganze Ausgabe dauert. Mit V3 kommen wir aber da nicht bei... :(

sidey79, 2017-12-05:
Ich habe hier nun mal eine Aufnahme meiner Anpassung. Es geht auch genau um dieses Protokoll.
Ich bin unsicher, was hier passiert ist. Die  Nachrichten können meiner Meinung Nach durch eine Sequenz 012 getrennt werden. 

```
MU;;P0=-11548;P1=5114;P2=-1844;P3=274;P4=-831;P5=625;P6=-460;D=12345634343456563434563434343434345656345634565656343434343434563434563456345634563434345634343456343434343456563434343434343434343434345634565634343434565656563434343434563456563434345634345634343434565634343434345656345634301234563434345656343456343434;
       P0=-11504;P1=5145;P2=-1816;P3=278;P4=-813;P5=648;P6=-465;D=34343456563456345656563434343434345634345634563456343434343434343434563456563434343456565656343434343456345656343434563434563434343456563434343434565634563430123456343434565634345634343434343456563456345656563434343434345634345634563456345634343456343434;
	   P0=-11524;P1=5170;P2=-1796;P3=275;P4=-826;P5=653;P6=-450;D=56343434343456563434343434343434343434345634565634365634343456343456343434345656343434343456563456343012345634343456563434563434343434345656345634565656343434343434563434563456345634563434345634343456343434343456563434343434343434343434345634565634343434;
	   P0=-11530;P1=5101;P2=-1794;P3=281;P4=-820;P5=626;P6=-459;D=56565656343434343456345656343434563434563434343456501234563434345656343456343434343434565634563456565634343434343456343456345634563456343434563434345634343434345656343434343434343434343434563456563434343456565656343434343456345656343434563434563434343456;
	   P0=-11530;P1=5101;P2=-1794;P3=271;P4=-813;P5=640;P6=-456;D=56343434343456563456343;CP=3;R=55;o;
```

Es sind aber nur zwei identische Wiederholungen enthalten (1 und 5), was mich ein wenig verwundert.
Vielleicht ist uns der Puffer hier voll gelaufen, das wäre eine Erklärung. Ich müsste mal messen, wie lange die Ausgabe der 1. Nachricht gedauert hat und gut der Puffer zu diesem Zeitpunkt gefüllt war.

@Ralf9 
Wie wäre das mit deiner MU Cut variante? Bis P0 würde hier ein buffermove gemacht oder?

Ralf9, 2017-12-05:
Ja bei P0 würde ein buffermove gemacht

sidey79, 2017-12-06:
Ich habe gestern noch die Info bekommen, dass die Wiederholungen nicht identisch gesendet werden.

Was das bedeutet weiss ich noch nicht :(

SebastianJuch, 2017-12-07:
Hallo,
dann melde ich mich mal kurz als der Tippgeber das die Wiederholungen nicht identisch sind :-)

Es ist so, ich habe die Puls folgen damals alle mit dem Oszi aufgenommen und gespeichert. habe diese dann händisch ausgewertet und angefangen zu decodieren.
CH5 Stop Signal bei meinem Handsender sieht bei einem Tastendruck folgendermaßen aus.

`

F | 0 | F | Q | F | 0 | 0 | 0 | 1 | F | F | 1 | 0 | 0 | F | F | 0 | F | 0 | F | 0 | F | 0 | F | 0 | 0 | F | Q | 0 | 0 | 0 | 0 | 0 | F | F | Q | 0 | F | 1 | Q | 0 | 0 | Q | Q | 0 | 0 | 1 | Q | 0 | 0 | F | Q | 1 | F | 1 | 0
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
F | 0 | F | Q | F | 0 | 0 | 0 | 1 | F | F | 1 | 0 | 0 | F | F | 0 | F | 0 | F | 0 | F | 0 | F | 0 | 0 | F | Q | 0 | 0 | 0 | 0 | 0 | F | F | Q | 0 | F | 1 | Q | 0 | 0 | Q | Q | 0 | 0 | 1 | Q | 0 | 0 | F | Q | 1 | F | 1 | 0
F | 0 | F | Q | F | 0 | 0 | 0 | 1 | F | F | 1 | 0 | 0 | F | F | 0 | F | 0 | F | 0 | F | 0 | F | 0 | 0 | F | Q | 0 | 0 | 0 | 0 | 0 | F | F | Q | 0 | F | 1 | Q | 0 | 0 | Q | Q | 0 | 0 | 1 | Q | 0 | 0 | F | Q | 1 | F | 1 | 0
F | 0 | F | Q | F | 0 | 0 | 0 | 1 | F | F | 1 | 0 | 0 | F | F | 0 | F | 0 | F | 0 | F | 0 | F | 0 | 0 | F | Q | 0 | 0 | 0 | 0 | 0 | F | F | Q | 0 | F | 1 | Q | 0 | 0 | Q | Q | 0 | 0 | 1 | Q | 0 | 0 | F | Q | 1 | Q | Q | F
F | 0 | F | Q | F | 0 | 0 | 0 | 1 | F | F | 1 | 0 | 0 | F | F | 0 | F | 0 | F | 0 | F | 0 | F | 0 | 0 | F | Q | 0 | 0 | 0 | 0 | 0 | F | F | Q | 0 | F | 1 | Q | 0 | 0 | Q | Q | 0 | 0 | 1 | Q | 0 | 0 | F | Q | 1 | Q | Q | F
F | 0 | F | Q | F | 0 | 0 | 0 | 1 | F | F | 1 | 0 | 0 | F | F | 0 | F | 0 | F | 0 | F | 0 | F | 0 | 0 | F | Q | 0 | 0 | 0 | 0 | 0 | F | F | Q | 0 | F | 1 | Q | 0 | 0 | Q | Q | 0 | 0 | 1 | Q | 0 | 0 | F | Q | 1 | Q | Q | F
F | 0 | F | Q | F | 0 | 0 | 0 | 1 | F | F | 1 | 0 | 0 | F | F | 0 | F | 0 | F | 0 | F | 0 | F | 0 | 0 | F | Q | 0 | 0 | 0 | 0 | 0 | F | F | Q | 0 | F | 1 | Q | 0 | 0 | Q | Q | 0 | 0 | 1 | Q | 0 | 0 | F | 1 | 1 | Q | Q | Q
`
Man sieht das die letzten 3 Stellen (6 Bits) variieren. Das ist bei allen Kanälen und Befehlen so.
Anbei noch die Auswertung als Excel Tabelle.

Gerne nehme ich das alles nochmal mit dem jetzt vorhandenen Logic Analyser auf um möglich Fehler auszuschließen.
[ACOMAX Handsender.xlsx](https://github.com/RFD-FHEM/SIGNALDuino/files/1539950/ACOMAX.Handsender.xlsx)

Gruß Sebastian

sidey79, 2017-12-08:
Ich fasse meine Sicht zusammen:

Die Nachricht hat 114 Bit
Die Nachricht wird 7 Mal gesendet, jedoch sind die Wiederholungen nicht identisch.

In der experimentellen Firmware waren manche Wiederholungen kürzer, was auf Datenverlust hinweist.
Wir müssten jetzt aussortieren, ob wir alle Wiederholungen brauchen oder ob uns eine ausreicht.

sidey79, 2017-12-16:
Ich probier noch mal, ob man die Nachrichten mit einer langen Pause nicht als MS ausgeben sollte.

Ansonsten zum Thema sehr lange Nachrichten habe ich mir ein paar Grundlegende Dinge überlegt. Ist aber vielleicht eher was für Version 4 :)

Sobald wir die minBitLength überschritten haben, könnten wir mit der Ausgabe (sofern es nicht Manchester ist) beginnen. Wir müssten die Ausgabe D= halt nur weiter nach Vorne ziehen.

Be Manchester Signalen, könnte man auch schon was ausgeben, wenn der Puffer komplett demoduliert wurde wurde, wir aber am Ende nichts manchester untypisches gefunden haben

SebastianJuch, 2017-12-18:
So, hier das versprochene Update von der heute eingespielten experimentellen Version von dir Sidney.

`2017.12.18 21:58:29 4: Funk1/msg READ: MC;LL=-798;LH=665;SL=-434;SH=295;D=49249;C=365;L=20;R=69;
2017.12.18 21:58:29 4: Funk1/msg READ: MC;LL=-796;LH=661;SL=-435;SH=296;D=49249;C=364;L=20;R=69;
2017.12.18 21:58:29 4: Funk1/msg READ: MC;LL=-792;LH=655;SL=-444;SH=303;D=49249;C=365;L=20;R=69;
2017.12.18 21:58:29 4: Funk1/msg READ: MU;P0=5222;P1=-1683;P2=291;P3=-454;P5=-807;P6=643;P7=-11546;D=25256325252563252525252563632525252525252525252525256325636325252525636363632525252525632563632525256325636325252563256325256325632525636325270125632525256363252563252525252525636325632563636325252525256325632563256325632563252525632525256325252525256363;CP=2;R=69;O;
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 49 -> quigg_gt9000 matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2017.12.18 21:58:29 4: Funk1: WS2000 Sensortyp 13 - ERROR typ to big
2017.12.18 21:58:29 4: Funk1: WS2000 Sensortyp 13 - ERROR typ to big
2017.12.18 21:58:29 4: Funk1: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 69 -> Hoermann matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 72 -> Siro shutter matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1/msg READ: MU;P0=288;P1=-801;P2=650;P3=-462;P4=-11436;P5=5220;P6=-1772;D=045601230101012323010123010101010101232301230123232301010101012301230123012301230123010101230101012301010101012323010101010101010101010101230123230101010123232323010101010123012323010101;CP=0;R=69;O;
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 49 -> quigg_gt9000 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 69 -> Hoermann matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 72 -> Siro shutter matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1/msg READ: MU;P0=638;P1=-815;P2=288;P3=-456;P4=-11484;P5=5224;P6=-1760;D=2456210321212103032121032121212121210303210321030303212121212103210321032103210321032121210321212103212121212103032121212121212121212121210321030321212121030303032121212121032103032121210321030321212103210321210321;CP=2;R=69;O;
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 49 -> quigg_gt9000 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 69 -> Hoermann matches, trying to demodulate
[... cut ...]
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 72 -> Siro shutter matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2017.12.18 21:58:30 4: Funk1/msg READ: MC;LL=-814;LH=655;SL=-444;SH=285;D=49249;C=366;L=20;R=71;
2017.12.18 21:58:30 4: Funk1/msg READ: MC;LL=-801;LH=647;SL=-447;SH=291;D=49249;C=364;L=20;R=71;`

sidey79, 2017-12-18:
Meine Idee hat offenbar nicht so richtig funktioniert. Hmm

SebastianJuch, 2017-12-18:
Was ist den damit?  Habe mich nochmal direkt vor dem Empfänger gestellt!!

```
2017.12.18 22:08:03 1: Logfile gelöscht
2017.12.18 22:08:12 4: Funk1/msg READ: MU;P0=-846;P1=246;P2=617;P3=-489;P4=-11280;P5=4980;P6=-1824;D=01010102323102310232323101010101010102310231023102310231010102310101023101010101023231010101010101010101010102310232310101010232323231010101010231023231010231010101010101010232310101010231010101010245610231010102323101023101010101010232310231023232310101;CP=1;R=101;O;
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 49 -> quigg_gt9000 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 69 -> Hoermann matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 72 -> Siro shutter matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1/msg READ: MU;P0=-849;P1=238;P2=585;P3=-507;P4=-11120;P5=5140;P6=-1824;D=01023231013102310231023102310101023101010231010101010232310101010101023102323101023101010101010101023231010101023101010101024561023101010232310102310101010101023231023102323231010101010101023102310231023102310101023101010231010101010232310101010101010101;CP=1;R=101;O;
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 37 -> Bresser 7009994 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: decoded matched MU Protocol id 40 dmsg u40#55110605900C20 length 56 RSSI = -23.5
2017.12.18 22:08:12 4: SIGNALduino_unknown incomming msg: u40#55110605900C20
2017.12.18 22:08:12 4: SIGNALduino_unknown rawData: 55110605900C20
2017.12.18 22:08:12 4: SIGNALduino_unknown Protocol: 40
2017.12.18 22:08:12 4: SIGNALduino_unknown converted to bits: 01010101000100010000011000000101100100000000110000100000
2017.12.18 22:08:12 4: Unknown, please report
2017.12.18 22:08:12 3: Funk1: Unknown code u40#55110605900C20, help me!
2017.12.18 22:08:12 4: Funk1: decoded matched MU Protocol id 40 dmsg u40#81AE02AA220C00 length 56 RSSI = -23.5
2017.12.18 22:08:12 4: SIGNALduino_unknown incomming msg: u40#81AE02AA220C00
2017.12.18 22:08:12 4: SIGNALduino_unknown rawData: 81AE02AA220C00
2017.12.18 22:08:12 4: SIGNALduino_unknown Protocol: 40
2017.12.18 22:08:12 4: SIGNALduino_unknown converted to bits: 10000001101011100000001010101010001000100000110000000000
2017.12.18 22:08:12 4: Unknown, please report
2017.12.18 22:08:12 3: Funk1: Unknown code u40#81AE02AA220C00, help me!
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 49 -> quigg_gt9000 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 69 -> Hoermann matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 72 -> Siro shutter matches, trying to demodulate
2017.12.18 22:08:12 4: Funk1: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2017.12.18 22:08:13 4: Funk1/msg READ: MU;P0=-874;P1=235;P2=581;P3=-520;P4=-11376;P5=5156;P6=-1684;D=01010101010232310231023232310101010101010231023102310231023101010231010102310101010102323101010101010101010101010231023231010101023232323101010101023102323101023101010101010101023231010101023101010101024561023101010232310102310101010101023231023102323231;CP=1;R=99;O;
2017.12.18 22:08:13 4: Funk1: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2017.12.18 22:08:13 4: Funk1: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2017.12.18 22:08:13 4: Funk1: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
[... cut ...]
2017.12.18 22:08:16 4: Funk1: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2017.12.18 22:08:16 4: Funk1: Fingerprint for MU Protocol id 69 -> Hoermann matches, trying to demodulate
2017.12.18 22:08:16 4: Funk1: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2017.12.18 22:08:16 4: Funk1: Fingerprint for MU Protocol id 72 -> Siro shutter matches, trying to demodulate
2017.12.18 22:08:16 4: Funk1: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate

```

sidey79, 2017-12-18:
@SebastianJuch  

Das liegt nicht an dir. Meine Idee, wie man das Signal erkenne könnte hat nicht funktioniert.
Gut hinterher ist man halt schlauer.

Bin halt noch unsicher, welcher Weg der insgesamt bessere ist.

SebastianJuch, 2017-12-18:
Achso, weil er ihn immer noch als MU nimmt und nicht wie angenommen als MS.
-.-
Schade Schade.

Kann man nicht den Sync der am Anfang gesendet wird irgendwie als Start definieren?

sidey79, 2017-12-18:
Naja, die Sequenz scheint irgendwie -11376, 5156, -1684;  zu sein.

Den 5156 zu erkennen, da er nur 1x im Puffer vorkommt ist etwas Risiko behaftet. Da besteht halt immer die Chance, dass man für andere Übertragung etwas ungünstiges einbaut.

Vielleicht kann man aber auf die 11ms Sendepause gehen. Das scheint mir doch durchaus ein valider Anhaltspunkt

SebastianJuch, 2017-12-18:
Wenn ich den Handsender mit einem LA auslese ergibt sich ein STAND von 5250 HIGH, 1750 LOW. Die Sendepause beträgt dann nach der ersten Aussendung 11500 LOW. 
Ich weiß nicht ob du die rc-switch für den Raspberry kennst, aber ich habe die Auswertung seines Codes damals in eine Excel Tabelle nachgebaut und er macht es so ähnlich wie du geschrieben hast. Das erste Telegramm rauscht einfach so durch und erst die Pause in Verbindung mit dem SYNC erzeugt das Start Kriterium für eine Nachricht.

Vielleicht kann man es ja so ähnlich machen.

Gruß Sebastian