RFD-FHEM/SIGNALDuino #81: Sendekommando optimieren

Ralf9, 2017-12-07:
wie hier
https://github.com/RFD-FHEM/RFFHEM/issues/138#issuecomment-349430894
aufgefallen ist, gibt es beim Sendekommando noch einiges zu optimieren und verbessern.
Damit müsste man es hinbekommen, daß es auch ohne msg_part funktioniert:
```
bool split_cmdpart(int16_t *startpos)
{
	int16_t endpos=0;
	endpos=cmdstring.indexOf(";",*startpos);     			 // search next   ";"

	if (endpos ==-1 || *startpos== -1) return false;
	msg_cmd0 = cmdstring.charAt(*startpos);
	msg_cmd1 = cmdstring.charAt(*startpos+1);
	if (msg_cmd0 == 'P') {
		if (cmdstring.charAt(*startpos+2) != '=') return false;
		startdata = *startpos+3;
	} else {
		if (msg_cmd1 != '=') return false;
		startdata = *startpos+2;
	}
	*startpos=endpos+1;    // Set startpos to endpos to extract next part
	return true;
}
```

sidey79, 2017-12-07:
Besser wir bauen die Stringklasse aus.

Ralf9, 2017-12-07:
Wie willst Du es ohne den cmdstring lösen?

sidey79, 2017-12-07:
Einfache char arrays und standard C Funktionen

Ralf9, 2017-12-07:
Dann musst Du Dich beim Sendekommando auf eine maximale Länge festlegen.
Das char array muß dann so groß definiert werden, daß auch sehr lange Sendekommandos reinpassen

sidey79, 2017-12-07:
vom Prinzip ja. Hast Du dir mal angesehen was die string bibliothek macht?

Ralf9, 2017-12-07:
nein, was macht sie?
Ich dachte bei z.B. reserve(40) wird fest 40 reserviert, wenn mehr benötigt wird, wird es dynamisch dazu allociert

sidey79, 2017-12-07:
Es wird dynamisch allokiert richtig. Wenn Du 40 reservierst und dann 50 brauchst, wird versucht 50 byte zu allokieren.Wenn die am Stück nicht frei sind... pech... 
Früher gab es dann Salat. Der ist vermutlich behoben, aber wenn der Speicher nicht am Stück ist, dann ist ende Gelände.

Ralf9, 2017-12-07:
Woran ist dann der Unterschied (außer daß mit strings das Handling einfacher ist) ob wir z.B. string.reserve(200) oder  char cmd[200] verwenden?
Ist 200 groß genug?

Ralf9, 2017-12-07:
mit
get sduino freeram
bekomme ich momentan ca 930, wie groß muß das freeram mindestens sein?

sidey79, 2017-12-07:
https://hackingmajenkoblog.wordpress.com/2016/02/04/the-evils-of-arduino-strings/

Ralf9, 2017-12-07:
Wie willst Du das 
.substring(3).toInt();
mit einem char array realisieren?

sidey79, 2017-12-08:
Mit atoi und und strstr kann man einen Teilstring in ein int umwandeln:

http://www.cplusplus.com/reference/cstdlib/atoi/
http://www.cplusplus.com/reference/cstring/strstr/

die cstring Lib bietet noch einige andere Optionen wie srncpy
http://www.cplusplus.com/reference/cstring/strncpy/?kw=strncpy

oder einfach ein strtok um einzelne Werte die über ein Trennzeichen abgetrennt wurden zu separieren:
http://www.cplusplus.com/reference/cstring/strtok/?kw=strtok

Sind ein paar Zeilen mehr, aber vielleicht auch langfristig viele Probleme weniger

Ralf9, 2017-12-08:
Ja, damit müsste es auch mit einem char array machbar sein. Den cmdstring auf ein char array umzubauen dürfte recht aufwändig werden.

Ralf9, 2017-12-10:
Ich habe in meiner Version das send_cmd überarbeitet.
Ich habe das msg_part entfernt und Syntaxprüfungen zugefügt.
Das cmdstring.reserve habe ich auf 240 erhöht.
Das umstellen von String zum char array war mir zu aufwändig.

sidey79, 2017-12-10:
240 ist ein Brocken.

Vielleicht braucht man den Puffer so nicht, wenn man die ganze serielle Anbindung etwas umbaut.

Den haben wir ja nur, um eine ganze Übertragung speichern zu können.

Ralf9, 2017-12-10:
Mit dem FS10 Sendekommando kommen wir über 200 
> P61#0000000000001000110001100011100011000101111PPPP0000000000001101110001100011100011000100101PPPP#R3

Damit haben wir immer noch über 700 freeram, ist dies zuwenig?
Wie groß sollte das freeram mindestens sein

elektron-bbs, 2017-12-17:
@Ralf9 
Ich habe deine Version V 3.3.2-dev_2017-12-10_Ralf9 mal getestet. Leider funktioniert das Senden nicht 100%ig:
[image]
So sollte es aussehen:
[image]

elektron-bbs, 2017-12-17:
Ich habe auch noch ein wenig mit der Version V 3.3.1-dev-mcfirtbitfix_2017-12-14 und den Variablen cmdstring.reserve und msg_part.reserve experimentiert.

1. Versuch:
Firmware unverändert
Neustart free RAM: 749
FS20 senden OK
FS10 senden OK
nach Senden free RAM: 562 (diff 187)

2. Versuch
	//cmdstring.reserve(40);
Neustart free RAM: 787
FS20 senden OK
nach Senden free RAM: 620 (diff 167)
FS10 senden OK
nach Senden free RAM: 562 (diff 58)

3. Versuch
	//msg_part.reserve(30);
Neustart free RAM: 789
FS20 senden OK
nach Senden free RAM: 622 (diff 167)
FS10 senden OK
nach Senden free RAM: 564 (diff 58)

4. Versuch
	msg_part.reserve(240);
Neustart free RAM: 789
FS20 senden OK
nach Senden free RAM: 622 (diff 167)
FS 10 nicht OK - am Beginn fehlen vorn 3 Bit
FS 10 nicht OK - bei der Wiederholung fehlen 26 Bit
nach Senden free RAM: 564 (diff 58)

elektron-bbs, 2017-12-17:
Außerdem ist mir aufgefallen, das der Watchdog beim Senden mehrerer Wiederholungen zuschlägt. Zwei Sekunden sind bei FS10 dann offensichtlich zu wenig.

Entweder sollte die Zeit für den WD gleich auf 8 S gesetzt werden:
//	wdt_enable(WDTO_2S);  	// Enable Watchdog
  wdt_enable(WDTO_8S);    // Enable Watchdog

oder wdt_reset() müsste in die Sendeschleife:

  for (uint8_t i = 0; i < repeats; i++) {
    for (uint8_t c = 0; c <= cmdNo; c++) {
      if (command[c].type == raw) {
        for (uint8_t rep = 0; rep < command[c].repeats; rep++) {
          wdt_reset();
          send_raw(command[c].datastart, command[c].dataend, command[c].buckets);
        }
      }
      //if (command[c].type == raw) { for (uint8_t rep = 0; rep < command[c].repeats; rep++) send_raw(command[c].datastart, command[c].dataend, command[c].buckets); }
      if (command[c].type == manchester) {
        for (uint8_t rep = 0; rep < command[c].repeats; rep++)send_mc(command[c].datastart, command[c].dataend, command[c].sendclock);
      }
      digitalLow(PIN_SEND);
    }
    if (extraDelay) delay(1);
  }

Diese Variante funktioniert übrigens nicht:

	for (uint8_t i=0;i<repeats;i++)
	{
    wdt_reset();
		for (uint8_t c=0;c<=cmdNo;c++)
		{
			if (command[c].type == raw) { for (uint8_t rep = 0; rep < command[c].repeats; rep++) send_raw(command[c].datastart, command[c].dataend, command[c].buckets); }
			if (command[c].type == manchester) { for (uint8_t rep = 0; rep < command[c].repeats; rep++)send_mc(command[c].datastart, command[c].dataend, command[c].sendclock); }
			digitalLow(PIN_SEND);
		}
		if (extraDelay) delay(1);
	}

Ich verstehe nicht so richtig, warum "repeats" eigentlich zweimal verwendet wird.

Ralf9, 2017-12-17:
> Ich habe deine Version V 3.3.2-dev_2017-12-10_Ralf9 mal getestet. Leider funktioniert das Senden nicht 100%ig:

Welches sendraw Kommando verwendest Du?

elektron-bbs, 2017-12-17:
Das sieht so aus:

2017.12.17 19:40:22 3: sduino868: FS20 set FS20_9c6400 on
2017.12.17 19:40:22 3: sduino868: sendmsg Preparing rawsend command for protocol=74, repeats=3, clock=400 bits=00000000000011001110000110010010000000000001000100001011100P
2017.12.17 19:40:22 3: sduino868/set: sending via SendMsg: SR;R=3;P0=600;P1=-600;P2=400;P3=-400;P4=-10000;D=23232323232323232323232301012323010101232323230101232301232301232323232323232323232323012323230123232323012301010123234;

sidey79, 2017-12-17:
@elektron-bbs 

Ist das dein Ernst, dass bei FS10 das Frequenzband 2 Sekunden lang belegt wird?
Wir können den Watchdog natürlich auf 8 Sekunden hoch setzten. Dem Zweck des Watchdog tut das nichts.

Aber es erscheint mir seltsam, dass das Frequenzband doch so lange belegt wird.

Zu dem Sende Kommando und dem repeat.
Ursprünglich war es möglich Sendekommandos zu verketten oder einzelne Sendebefehle abzusetzen.

Verketten geht so:

`SC;SR;R=3;P0=1000;P1=-2000;D=01;SR;R=3;P0=-20000;D=0;`

Das war aber ziemlich eingeschränkt, denn das Resultat war, dass 3x der 1. Sendebefehl und dann 3x der 2. Sendebefehl abgearbeitet wurde.

Ich habe den Sendebefehl daher erweitert und auch eine Wiederholungs Angabe für die Verkettung ermöglicht:

`SC;R=2;SR;R=3;P0=1000;P1=-2000;D=01;SR;R=3;P0=-20000;D=0;

Damit hat man nun die Chance die Wiederholung für die ganze Kette aber auch für jedes Sendekommando zu definieren.
Ich glaube ich wurde dazu durch einen deiner Beiträge inspiriert. :-)

In dem Beispiel, wird die Kette nun 2x abgearbeitet. Und die Kette besteht 3x aus 1000,-2000 und wird von einer langen Pause 3x-20000 gefolgt.

Damit ist es viel einfacher möglich, sogenannte Preamble Sequenzen zu senden. Anstatt jeden Puls einzeln in einem Sendebefehl zu übergeben kann man diesen auch in einen Sendebefehl mit einer Wiederholungsangabe stecken.

Es kann sein, dass da bei meiner Implementierung ein Fehler ist. Ralf hatte sich da auch schon mal geäußert dass da was nicht funktioniert. Für genauere Hinweise, was nicht funktioniert, wäre ich dankbar.

elektron-bbs, 2017-12-17:
Das dauert nur solange, wenn viele Dimm-Befehle abgesetzt werden sollen, oder wie von Ralf gewünscht, mehrere Wiederholungen. Bei nur einem einfachen on/off werden nur knapp 280 mSek benötigt.

Verstehe ich das richtig, das ich bei SendCombined den Sendebefehl (D=..) "raw" übergeben muss, also nicht nur Bits, die gesendet werden sollen?
Muss ich für die erste und zweite Message jeweils P0 bis Px übergeben?
Das komplette Kommando würde dann z.B. bei FS10 etwa so aussehen:

SC;R=2;SR;R=1;P0=400;P1=-800;P2=-400;P3=-32400;D=020202020202020202020202010102020201010201010102020201010102020201020201020102010202013;SR;R=2;P0=400;P1=-800;P2=-400;P3=-32400;D=020202020202020202020202010102020201010201010102020201010102020201020201020102010202013

Macht zusammen 266 Byte :-(

sidey79, 2017-12-17:
@elektron-bbs 

Das ist bei dem SR (Send Raw) Befehl immer so, dass Du die Signaldaten übergibst und keine Bits.
Nur bei Manchester Encoding, kannst Du Bits übergeben.

Für jeden Send Raw Befehl musst Du die Pegelzeiten separat übergeben. Es ist ein eigenständiges Sendekommando. Da bei deinem Beispiel die Signaldaten identisch sind, brauchst Du das SendCombined nicht. Du kannst einfach eine Wiederholung des SendRaw Befehls nutzen.

elektron-bbs, 2017-12-17:
Der Einfachheit halber habe ich zweimal die gleichen Signaldaten eingefügt. Die sind normalerweise natürlich bei FS10 verschieden.

Einen Haken hat die Sache noch: Ich müsste in die 10_FS10.pm die Puls-/Pausezeiten fest "einmeißeln". Oder kann ich die irgendwie aus der 00_SIGNALduino.pm auslesen?

sidey79, 2017-12-17:
@elektron-bbs 

Wenn ich alles richtig verstanden habe, dann musst Du mehrere unterschiedliche Nachrichten senden. Richtig?

Prinzipiell müsste man auf die Daten aus dem Protokoll zugreifen können.

elektron-bbs, 2017-12-17:
Ja, bei FS10 werden immer 2 verschiedene Nachrichten im Abstand von 200 mSek gesendet. Die 2. Nachricht wird je nach Anzahl Dimmstufen wiederholt, auch wieder im Abstand von 200 mSek.

> Prinzipiell müsste man auf die Daten aus dem Protokoll zugreifen können.

Tja, aber wie? Mir gefiel es schon nicht besonders, die Protokoll-Id dort festzulegen.

sidey79, 2017-12-17:
Hmm, Theorie an:

Es gibt eine Funktion, die gibt Werte aus der Protokoll Liste zurück:
```

#=item SIGNALduino_getProtoProp()
#This functons, will return a value from the Protocolist and check if it is defined optional you can specify a optional default value that will be reurned
# 
# returns "" if the var is not defined
# =cut
#  $id, $propertyname,

sub SIGNALduino_getProtoProp
	my ($id,$propNameLst,$default) = @_;
	return $ProtocolListSIGNALduino{$id}{$propNameLst} if defined($ProtocolListSIGNALduino{$id}{$propNameLst});
	return $default; # Will return undef if $default is not provided
	#return undef;
}
```

Die Funktion ist vorhanden, ich habe es aber nicht ausprobiert, diese von einem anderen Modul aus zu verwenden. Das sollte aber funktionieren.
Ob man auf die Variable $ProtocolListSIGNALduino so einfach zugreifen kann weiss ich nicht. Eigentlich sollte das gehen. Eventuell muss man halt noch was mit dem Exporter machen oder wir bohren die Funktion SIGNALduino_getProtoProp etwas auf und übergeben ihr auch den hash vom IOdev.
Im Hash des iodev legen wir dann eine Referenz auf die ProtocolListSIGNALduino.
Klingt umständlich, sollte aber mit wenig Aufwand umsetzbar sein, wenns nicht direkt so funktioniert.

sidey79, 2017-12-18:
Die 200 msek Pause sind ja etwas blöde. Hmm. Hast Du es mal mit einer kürzeren Pause probiert?

elektron-bbs, 2017-12-18:
Habe ich nicht probiert. Der Hersteller wird sich schon etwas dabei gedacht haben. Die Pause stört doch keinen. Selbst wenn der SIGNALduino mal 2 Sekunden beschäftigt ist, hat er immer noch 86398 am Tag Zeit für andere Aufgaben :-)

elektron-bbs, 2017-12-18:
> Es kann sein, dass da bei meiner Implementierung ein Fehler ist. Ralf hatte sich da auch schon mal geäußert dass da was nicht funktioniert. Für genauere Hinweise, was nicht funktioniert, wäre ich dankbar.

Wie muss ich denn den Sendebefehl für SC denn jetzt absetzen?
Probiert habe ich:
IOWrite($hash, 'sendMsg', $newmsg);
IOWrite($hash, 'sendRaw', $newmsg);
IOWrite($hash, 'sendraw', $newmsg);
SIGNALduino_AddSendQueue($hash,$newmsg);
Brachte aber alles nichts.

Ralf9, 2017-12-18:
@elektron-bbs 
hast Du zum Testen ein FS10_DI oder FS10_HD?

Ich habe mal ein wenig getestet, dabei ist mir aufgefallen, daß bei der FS10 ST zum schalten nur das erste Kommando notwendig ist.
Ist dies bei den anderen FS10 Steckdosen genauso?
Demnach ist das zweite Kommando nur beim Dimmer notwendig.

sidey79, 2017-12-18:
@elektron-bbs 

Also die 200 ms Pause kann man vermutlich verschmerzen. Ich habe halt nur überlegt ob man die 200ms genau einhalten muss.
Die delayMicros Funktion ist dafür überhaupt nicht gedacht, so lange Pausen zu machen. 
Hätte ja sein können, dass es nicht so genau sein muss und man einfach das IOWrite um ca.200 ms verzögert und dadurch die Wiederholungen realisiert.

elektron-bbs, 2017-12-18:
@Ralf9 
Dimmer stehen mir leider nicht zur Verfügung. Ich halte mich nur daran, wie meine Fernbedienungen und der FS10 ZE senden. Warum sollte ich es anders als der Hersteller machen?

elektron-bbs, 2017-12-18:
@sidey79 
Man muss die Pause sicher nicht auf die Mikrosekunde genau einhalten. Es funktioniert ja jetzt auch wunderprächtig. Ich habe jetzt mit folgendem das SC getestet:

$newmsg ="SC;R=2;SR;R=1;P0=400;P1=-800;P2=-400;P3=-32400;D=02020202020202020202020201010202020101020101010202020101010202020102020201010101020101333333;SR;R=3;P0=400;P1=-800;P2=-400;P3=-32400;D=02020202020202020202020201020201020101020101010202020101010202020102020201010102010101333333;";
SIGNALduino_AddSendQueue($io,$newmsg);

Der 1. Teil wird einmal gesendet, der 2. dreimal und das ganze einmal wiederholt - insgesamt 8 Nachrichten, genau wie gewollt. Die Pausen stimmen auch. Der freie Speicher des SIGNALduino ist damit allerdings nochmal um 50 Byte kleiner geworden. In der Firmware habe ich noch die beiden Strings ohne ".reserve" drin und den Watchdog auf 8 Sekunden.

Mir gefällt nur nicht so richtig, das Sendekommando in der FS10.pm zusammen zu basteln. Besser aufgehoben wäre das meines Erachtens in der SIGNALduino.pm. Man müsste ja nur "sendMsg" um SC erweitern.

Ralf9, 2017-12-18:
Ich habe mal bei mir in die FS10_Set eine Abfrage eingebaut, damit, wenn kein Dimmer verwendet wird, nur das erste Komando gesendet wird.
Ich kann damit zwei FS10 ST zuverlässig schalten. Hat dies irgendwelche Nachteile die ich übersehen habe?
```
  $iNum = 1 if ($modelType ne "dimmer" );
  for my $i (1..$iNum) {
```
@elektron-bbs 
Woher weist Du, daß dies so mit dem SC Kommando, mit dem die gesammte Dimmnachricht mehrmals gesendet wird, so funktioniert?
Nach meinem Verständnis müsste das erste Kommando mehrmal gesendet werden und das zweite so oft wie es für das Dimmen notwendig ist.

elektron-bbs, 2017-12-19:
Weil in der Bedienungsanleitung zur FS10 S8 steht: "Die Betätigung erfolgt je nach Erfordernis kurz (Schalten) oder länger als 0,4 s (z.B. Dimmen)." Oszillogramme der FB habe ich aufgenommen und weiß dadurch, was diese sendet.

Schalten:
1. Nachricht
Pause 200 mSek
2. Nachricht

Dimmen:
1. Nachricht
Pause 200 mSek
2. Nachricht
Pause 200 mSek
2. Nachricht
Pause 200 mSek
2. Nachricht
... u.s.w.

1. und 2. Nachricht unterscheiden sich jeweils durch einen anderen Befehl und Prüfsumme.
Wir hatten das eigentlich schon mal durch in [](https://github.com/RFD-FHEM/RFFHEM/issues/138)

Ralf9, 2017-12-19:
Bei meiner FS10 S8 konnte ich keine Wiederholungen feststellen, 
Konntest Du bei Deinen Fernbedienungen und der FS10 ZE beim Schalten und beim Dimmen Wiederholungen feststellen?

Ralf9, 2017-12-19:
Da es beim Dimmer keine Wiederholung gibt, könnte man dies, wie von Sidey auch schon vorgeschlagen mit einem `InternalTimer(gettimeofday() + 0.2, ...` lösen.
Die erste Nachricht normal senden und dann die 2. und folgende Nachrichten mit dem InternalTimer verzögern.

Beim Schalten ist mir aufgefallen, daß es auch funktioniert, wenn man nur die 1.Nachricht sendet.
Das Schalten hat auch funktioniert, wenn ich bei dieser Nachricht ein Repeat (z.B. #R3) verwendet habe.
Hat dies, wenn kein Dimmer verwendet wird, irgendwelche Nachteile?

Beim Schalten mit Dimmer fallen mir die folgenden Varianten ein:
a)
```
1.Nachricht ggf mit Repeat
200ms Pause
2. Nachricht  ggf mit Repeat
```
b)
```
  ( 1.Nachricht,  200ms Pause,  2. Nachricht,  200ms Pause ) #Repeat n
```

Hast Du getestet welche Variante funktioniert.
Wir haben dabei das Problem, daß wir nicht testen können, ob und welche Variante beim Dimmer fuktioniert.

elektron-bbs, 2017-12-19:
> Bei meiner FS10 S8 konnte ich keine Wiederholungen feststellen

Wieso jetzt nicht mehr? Am 19.03.2017 hattest du die Wiederholungen selbst noch dokumentiert:
[https://github.com/RFD-FHEM/RFFHEM/issues/138#issuecomment-287636086](url)

> Ich habe auch eine FS10 S8
 Dies erhalte ich wenn ich eine Taste drücke.
 Habt ihr eine Protokollbeschreibung gefunden?

Taste1 off
 P61#00094638C78
 P61#000C4638C60
 Taste1 on
 P61#000DC638C74
 P61#0008C638C5C

Natürlich kann ich bei meiner Fernbedienung und dem Timer die Wiederholungen sehen. Ich hätte hier z.B. einen Auszug aus dem Log vom PC-Wetter Sensor-Testempfaenger:

2017/12/19 16:53:20: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 0 Befehl: 1
2017/12/19 16:53:21: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 1 Befehl: 1
2017/12/19 17:01:27: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 0 Befehl: 0
2017/12/19 17:01:27: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 1 Befehl: 0
2017/12/19 17:20:22: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 0 Befehl: 1
2017/12/19 17:20:22: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 1 Befehl: 1
2017/12/19 18:53:01: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 0 Befehl: 0
2017/12/19 18:53:01: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 1 Befehl: 0
2017/12/19 20:24:38: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 0 Befehl: 1
2017/12/19 20:24:38: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 1 Befehl: 1
2017/12/19 20:39:59: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 0 Befehl: 1
2017/12/19 20:40:00: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 2 Befehl: 1
2017/12/19 20:40:00: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 2 Befehl: 1
2017/12/19 20:40:00: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 2 Befehl: 1
2017/12/19 20:40:01: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 2 Befehl: 1
2017/12/19 20:40:01: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 2 Befehl: 1
2017/12/19 20:40:01: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 2 Befehl: 1
2017/12/19 20:40:01: remote_control: HC: 1 Typ: 0 Gr: 1 Adr: 1 Wh: 2 Befehl: 1

Die letzten Befehle ab 20:39:59 kommen beim Festhalten einer Taste (Dimmen).

Das einfache Schalten funktioniert bei mir auch meistens mit der 1. Nachricht. Nachteile haben die Wiederholungen keine. Es erhöht nur die Zuverlässigkeit.
Das Senden funktioniert zuverlässig mit der neu eingebauten Pause und die benötige ich sowieso schon für die FS20. Ob nun 10 oder 200 mSek Pause verwendet werden, so werden die Zeiten exakt eingehalten. In Perl realisiert dürfte das nicht so gut funktionieren, zumal der SIGNALduino in der Pause schon wieder auf Empfang geht.
Zum Dimmen ist eigentlich wirklich das SC ideal geeignet. Ich würde jetzt ein neues Kommando "sendMsgCombined" in die SIGNALduino_Set einbauen. Das "sendMsg" umzubauen für alle Varianten scheint mir zu umständlich.
Ich hänge mal noch meine aktuelle FS10.pm mit an. Da wird das Dimmen noch mit zwei getrennten sendMsg ausgeführt. Das hat allerdings den Nachteil, das eine zusätzliche Verzögerung von ca. 55 mSek die Pause verlängert. Dies dürfte auf irgendwelche Programmlaufzeiten zurückzuführen sein und ist dann sicher noch unterschiedlich je nach verwendeter Hardware.
[10_FS10.zip](https://github.com/RFD-FHEM/SIGNALDuino/files/1573075/10_FS10.zip)

Ralf9, 2017-12-19:
> Wieso jetzt nicht mehr? Am 19.03.2017 hattest du die Wiederholungen selbst noch dokumentiert:

Dies sind keine Wiederholungen dies sind Nachricht 1 + 2.

> Natürlich kann ich bei meiner Fernbedienung und dem Timer die Wiederholungen sehen. Ich hätte hier z.B. einen Auszug aus dem Log vom PC-Wetter Sensor-Testempfaenger

Bei Deinem Log, sind dies Wiederholungen der Variante a?

```
1.Nachricht ggf mit Repeat
200ms Pause
2. Nachricht  ggf mit Repeat
```

Mit dem sendMsg lässt sich dies momentan nur mit einzelnen sendMsg lösen.
Ohne einen Dimmer wissen wir nicht wie genau die 200ms sein müssen.
Ich würde erstmal eine einfache Variante einbauen. Wenn sich jemand meldet mit einen Dimmer können wir es testen und ggf verbessern

**Fall1a schalten mit Repeat:**
sendMsg(..P61#xxxxxxxP#Rn)        für Nachricht 1
sendMsg(..P61#PPPP#R1)             für die 200ms Pause
sendMsg(..P61#xxxxxxxP#Rn)        für Nachricht 2

**Fall1b schalten ohne Repeat:**
sendMsg(..P61#xxxxxxxPPPPxxxxxxx#R1

**Fall2 dimmen:**
sendMsg(..P61#xxxxxxxP#Rn)        für Nachricht 1
sendMsg(..P61#PPPP#R1)             für die 200ms Pause
sendMsg(..P61#xxxxxxxPPPP#Rdimm)        für Nachricht 2

Ralf9, 2017-12-23:
@sidey79 
Das Sendekommando lässt sicht recht einfach deutlich verkürzen
ich habe als Beispiel mal Dein OSV2 Sensor genommen
```
SR;P2=-1034;P3=922;P4=-547;P5=425;P6=-9356;D=32323232323232323232323232323232323452543452543452543232345232543234523254345232323232543452323232323232323254345232323254345254323452325432345232543452323232323254345232543232345232323232323232543236323232323232323232323232323232345254345254345254323234523254323452325434523232323254345232323232323232325434523232325434525432345232543234523254345232323232325434523254323234523232323232323254323;
SR;a=23;P2=-1034;P3=922;P4=-547;P5=425;P6=-9356;D=3aaaaaaaaaaaaaaaaa452543452543452543aa45a2543a45a254345aaaa254345aaaaaaaa254345aaa2543452543a45a2543a45a254345aaaaa254345a2543aa45aaaaaaa2543a63aaaaaaaaaaaaaaa452543452543452543aa45a2543a45a254345aaaa254345aaaaaaaa254345aaa2543452543a45a2543a45a254345aaaaa254345a2543aa45aaaaaaa2543a;
SR;a=23;b=45;P2=-1034;P3=922;P4=-547;P5=425;P6=-9356;D=3aaaaaaaaaaaaaaaaab2543b2543b2543aaba2543aba2543baaaa2543baaaaaaaa2543baaa2543b2543aba2543aba2543baaaaa2543ba2543aabaaaaaaa2543a63aaaaaaaaaaaaaaab2543b2543b2543aaba2543aba2543baaaa2543baaaaaaaa2543baaa2543b2543aba2543aba2543baaaaa2543ba2543aabaaaaaaa2543a;
```
Die 1.Zeile (orginal) hat ein Länge von 442
Bei der 2. Zeile habe ich 23 durch a ersetzt, die Länge ist nun 335
Bei der 3. Zeile habe ich 45 durch b ersetzt, die Länge ist nun 312

Hier ist als weiteres Beispiel eine MU-Nachricht:
```
SR;P0=-420;P2=498;P3=-1032;P4=1467;D=0232323232323234323434343232323234343434343232323432323434323432343232343434343434343432343234343434323434343434343434343234343434343434323434343434343232343434323234323432;
SR;a=23;P0=-420;P2=498;P3=-1032;P4=1467;D=0aaaaaaa43a434343aaaa4343434343aaa43aa4343a43a43aa4343434343434343a43a43434343a434343434343434343a43434343434343a434343434343aa434343aa43a432;
SR;a=23;b=43;P0=-420;P2=498;P3=-1032;P4=1467;D=0aaaaaaababbbaaaabbbbbaaabaabbababaabbbbbbbbababbbbabbbbbbbbbabbbbbbbabbbbbbaabbbaabab2;
```
Die 1.Zeile (orginal) hat ein Länge von 211
Bei der 2. Zeile habe ich 23 durch a ersetzt, die Länge ist nun 185
Bei der 3. Zeile habe ich 43 durch b ersetzt, die Länge ist nun 136

Dies in der send_raw Routine zu berücksichtigen dürfte kein großer Aufwand sein

Ralf9, 2017-12-23:
> Du hast mit SR gesendet, interessant ist halt was im Signalpuffer wirklich auf der Empfangsseite steht.
> Senden und Empfangen ist meistens nicht identisch. Ich habe das ganze getestet in dem ich die beschrieben Signale in den Puffer geschoben habe.

Daß das mit dem SR senden nicht so gut ist, habe ich inzwischen auch erkannt.
Es müsste doch auch funktionieren, wenn ich in der send_raw Routine anstatt zu senden, die Pulse direkt dem Decoder übergebe.
`#define SENDTODECODER`
```
void send_raw(const uint8_t startpos,const uint16_t endpos,const int16_t *buckets, String *source=&cmdstring)
{
	uint8_t index=0;
	unsigned long stoptime=micros();
	bool isLow;
	int16_t dur;
	for (uint16_t i=startpos;i<=endpos;i++ )
	{
		//MSG_PRINT(cmdstring.substring(i,i+1));
		index = source->charAt(i) - '0';
		//MSG_PRINT(index);
		isLow=buckets[index] >> 15;
		dur = abs(buckets[index]); 		//isLow ? dur = abs(buckets[index]) : dur = abs(buckets[index]);
#ifndef SENDTODECODER
		while (stoptime > micros()){
			;
		}
		isLow ? digitalLow(PIN_SEND): digitalHigh(PIN_SEND);
		stoptime+=dur;
#else
		if (isLow) dur = -dur;
		musterDec.decode(&dur);
#endif
	}
#ifndef SENDTODECODER
	while (stoptime > micros()){
		;
	}
#endif
}
```

Darf ich die #ifdef auch verschachteln?
```
#ifndef SENDTODECODER
...
#ifdef CMP_CC1101
...
#endif
#endif
```

sidey79, 2017-12-23:
Verschachteln von ifdef geht.
Allerdings lässt es sich einfacher analysieren, wenn man dafür einen separaten Sketch nimmt.

Ich habe das in dem SignalDecoderTester eingebaut, dass aus einem String das Signal erzeugt wird.
Ich habe auch eine 2. Variante, es direkt in den Puffer zu schreiben.

sidey79, 2017-12-23:
Verschachteln von ifdef geht.
Allerdings lässt es sich einfacher analysieren, wenn man dafür einen separaten Sketch nimmt.

Ich habe das in dem SignalDecoderTester eingebaut, dass aus einem String das Signal erzeugt wird.
Ich habe auch eine 2. Variante, es direkt in den Puffer zu schreiben.

Zu dem Reduzieren der Nachricht könnte man auch dir Message compress vom senden her verwenden. Das halbiert halt maximal.

Ralf9, 2017-12-23:
Es sollte lesbar bleiben und ich möchte das reduzierte Sendekommando auch über die serielle Console und über Telnet senden können.
Wenn ich anstatt a und b, z und o nehme sind in der 00_signalduino sendMsg Routine nur sehr geringe Anpassungen nötig.
SendMsg P61#0000000000001100010010100011100011000101111#R1
ergibt
`SR;R=1;o=02;z=01;P0=400;P1=-400;P2=-800;D=zzzzzzzzzzzzoozzzozzozozzzooozzzoozzzozoooo;`

Bei der ProtocolID 17 (itv3) reduzieren sich die Daten um den Faktor 4
sendMsg=P17#01010110010100000101111110010001#R6
ergibt
`SR;R=6;o=0302;z=0203;P0=250;P1=-2500;P2=-250;P3=-1250;D=01zozozoozzozozzzzzozoooooozzozzzo;`

sidey79, 2017-12-23:
Vielleicht sollte man noch mal drüber nachdenken. Das ist ja eine Aufgabe für eine klassische Komprimierung. Da gibt es ja schon viele Methoden

elektron-bbs, 2017-12-23:
@Ralf9 
Warum willst du die Bits eigentlich erst in Buchstaben wandeln und dann im Arduino wieder zurück?
Sinnvoller wäre dann vielleicht:
SR;R=1;0=400,-400;1=400,-800;D=0000000000001100010010100011100011000101111
Das spart auch schon wieder 11 Byte,
Oder die Daten in Hex und eine Bitanzahl dazu, weil wir es ja hier z.B. mit 43 Bits zu tun haben.
Allerdings würde ich dem kleinen Arduino nicht zu viel Arbeit aufhalsen, nur um ein paar Byte RAM zu sparen.

sidey79, 2017-12-23:
@elektron-bbs   @Ralf9 

Über das Speichern von Sequenzen anstellen von einzelnen Pulsen hatte ich schon mal ganz zu Beginn des Projektes nachgedacht.
Damals hauptsächlich von der Empfangsseite her. Ein Muster kann ja aus mehr als einem Puls bestehen.

Ich habe es damals allerdings verworfen, da ich Probleme hatte, die "richtigen" Pulse als Sequenz zu ermitteln. 
Beim Senden ist das vermutlich wesentlich einfacher zu handhaben.

Mit üblichen Kompressionsmethoden hatte ich mich auch schon mal beschäftigt. 

Man könnte das am Beispiel von udo wie folgt realisieren;
Übergeben von Sequenzen anstatt Pulsen mit Sx=  
`SR;R=1;S0=400,-400;S1=400,-800;D=0000000000001100010010100011100011000101111`
Die einfachste Variante einer Komprimierung wäre eine Anzahl mit der Sequenz zu übermitteln

`SR;R=1;S0=400,-400;S1=400,-800;D=0*12.1100010010100011100011000101111`

Leider benötigt man in diesem Fall ein Endezeichen. Exemplarisch mal der "." nach der Anzahl von 12 mal Sequenz 0. Diese Form nutz halt erst ab einer Wiederholung von 6x dem gleichen Puls in Folge. Das passiert halt nicht so oft.

Ralf9, 2017-12-23:
Beim ITv3 ist Deine Idee länger:
```
SR;R=6;o=0302;z=0203;P0=250;P1=-2500;P2=-250;P3=-1250;D=01zozozoozzozozzzzzozoooooozzozzzo;
SR;R=6;0=250,-250,250,-1250;1=250,-1250,250,-250;P2=250;P3=-2500;D=2301010110010100000101111110010001;
```
Ich möchte mit überschaubaren Aufwand das Sendekommando reduzieren.

Ralf9, 2017-12-23:
Die Sendereduzierung sollte auch funktionieren, wenn ich eine Nachricht empfange und dann wieder senden will.
z.B.
`MU;P0=32001;P1=-1033;P2=506;P3=1465;D=0121212121212121213121313131212121213131313131212121312131313121312131213131213131313131312131313131313131212121313131313131213131313131313121313131313131213131213121313131312;CP=2;R=39;`
wenn ich dann dies sende
`get sduino raw SR;Z=12;O=13;P0=32001;P1=-1033;P2=506;P3=1465;D=012121212...`
dann soll in der 00_Signalduino die Z und O nach z und o gewandelt und in den Daten eingetragen werden
`SR;z=12;o=13;P0=32001;P1=-1033;P2=506;P3=1465;D=0zzzzzzzzozo...`

sidey79, 2017-12-23:
Was soll Z oder O bedeuten?

Ralf9, 2017-12-23:
z=zero
o=one
Wenn es beim get sduino raw das z und o gross geschrieben wird, weiß die Routine, daß sie vor dem Senden das z und o noch in die Daten eintragen muss

Ralf9, 2017-12-23:
ich habe noch keine so richtige Idee wie man das am besten in die send_raw Routine einbaut?
Eine Möglichkeit wäre die startpos und endpos von o und z zu übergeben
`void send_raw(const uint8_t zStartpos, const uint8_t zEndpos, const uint8_t oStartpos, const uint8_t oEndpos, const uint8_t startpos,const uint16_t endpos,const int16_t *buckets, String *source=&cmdstring)`

sidey79, 2017-12-23:
Du bist schon beim Implementieren, ich bin noch am überlegen was überhaupt implementiert werden sollte.

Mit Z und O bin ich nicht so richtig zufrieden. Was genau soll one oder zero bei einer Signalfolge bedeuten?  Wir sind hier doch auf einer Signalebene und können Sequenzen oder Pulse übergeben.

Das eine Sequenz  1 aus den Pulsen 1 und 2 bestehen kann würde ich verstehen. Auf zwei Pulse ist das ganze ja aber nicht begrenzt oder?

Ralf9, 2017-12-23:
in der 00_Signalduino sendraw habe es bei mir schon eingebaut, waren nur ein paar sehr kleine Anpassungen, da ich die Bezeichnungen für 0 und 1 aus der Protokolldefinition verwendet habe.
```
		foreach my $item (qw(sync start zero one float pause))
		{
		    #print ("item= $item \n");
		    next if (!exists($ProtocolListSIGNALduino{$protocol}{$item}));
		    
			foreach my $p (@{$ProtocolListSIGNALduino{$protocol}{$item}})
			{
			    #print (" p = $p \n");
			    
			    if (!exists($patternHash{$p}))
				{
					$patternHash{$p}=$cnt;
					$pattern.="P".$patternHash{$p}."=".$p*$clock.";";
					$cnt++;
				}
		    	$signalHash{$item}.=$patternHash{$p};
			   	#print (" signalHash{$item} = $signalHash{$item} \n");
			}
			if ($item eq 'one' || $item eq 'zero') {
				$pattern = substr($item,0,1) . '=' . $signalHash{$item} . ';' . $pattern;
			}
		}
		my @bits = split("", $data);
	
		my %bitconv = (1=>"one", 0=>"zero", 'D'=> "float", 'P'=> "pause");
		my $SignalData="D=";
		
		$SignalData.=$signalHash{sync} if (exists($signalHash{sync}));
		$SignalData.=$signalHash{start} if (exists($signalHash{start}));
		foreach my $bit (@bits)
		{
			next if (!exists($bitconv{$bit}));
			#Log3 $name, 5, "encoding $bit";
			if ($bit eq '0' || $bit eq '1') {
				$SignalData.= substr($bitconv{$bit},0,1);
			} else {
				$SignalData.=$signalHash{$bitconv{$bit}}; ## Add the signal to our data string
			}
		}
		$sendData = "SR;R=$repeats;$pattern$SignalData;$frequency";

```

Beim Arduino habe ich noch keine so richtige Idee wie man es da am Besten einbauen kann.

Ralf9, 2017-12-23:
0 und 1 kann ich nicht verwenden, da es schon von P0 und P1 verwendet wird

sidey79, 2017-12-23:
Also bislang stehen in D= die Signalpulse. Jetz möchtest Du Seqenzen in D= ablegen.

Soll in D= nur Sequenzen oder Sequenzen und Pulse gleichzeitig hinterlegt werden können, oder reicht es aus Sequenzen oder Pulse dort zu hinterlegen?

Ralf9, 2017-12-23:
Ich habe gedacht für 0 und 1 Sequenzen und für den Rest Pulse (z.b. sync, start oder pause)

Ralf9, 2017-12-24:
Sehe ich das richtig, daß beim send combined das SC am Anfang nur benötigt wird, wenn ein combined Repeat größer 1 benötigt wird.
Mir ist nicht klar zu was und in welchen Fällen das extraDelay benötigt wird.

[5 more replies]

closed 2020-04-12 by sidey79