RFD-FHEM/SIGNALDuino #82: Erkennung von MC-Nachrichten mit Sync verbessern

Ralf9, 2017-12-08:
hier #75 ist aufgefallen, daß es beim Oregon Scientific V1 Protocol Probleme beim Erkennen gibt.

Ich habe mir mal die Oregon Protokollbeschreibung angesehen:
```
Version 1.0 sensors have a simpler format as shown in figure 2 below.
1. The preamble contains twelve “1” bits.
2. The sync section consists of a long off period (4.2msec), a long RF pulse
(5.7msec) and another long off period (around 5msec).
3. The first data sample point (clock edge) is not always marked by an RF
transition and must be measured from the end of the long sync pulse.
```
Dies ist der Sync:
P3=-4320;P4=5680;P5=-6752;D=345
je nachdem ob das erste Datenbit 0 oder 1 ist, ist P5 5000-5400 oder 6480-6800

**Beim Somfy Protokoll gibt es wahrscheinlich ähnliche Probleme.**
Die Tastendrücke der Fernbedienung werden nicht zuverlässig erkannt, da die Nachricht ab und zu um 1 Bit verschoben ist.

ich habe mir das somfy-rts-protocol mal angeschaut.
https://pushstack.wordpress.com/somfy-rts-protocol/

Das Problem beim somfy ist, daß es im Gegensatz zu den anderen manchester Protokollen keine Preamble gibt.
Nach dem Software sync kommen direkt die Daten.
Evtl werden die "1/2 symbol width low(~604 us)" vom Software sync von der Firmware ab und zu als Daten erkannt.

Ralf9, 2017-12-08:
Die Erkennung müsste eigentlich zuverlässiger werden, wenn wir ein paar Syncpulse vor dem mc_start mit ausgeben z.B.
MC;P4=5712;P5=-6752;P1=1673;LL=-2618;LH=3139;SL=-1260;SH=1676;D=A7D3FF8C;C=1448;L=31;R=240;

In der 00_Signalduino können dann die Syncpulse verarbeitet werden und ggf der beginn der MC-Nachricht angepasst werden

Ich habe es mir so gedacht:
```

isManchester()
...
	mc_sync_count = 0;
	if (mc_start_found && pdec->mstart > 0)
	{
		for (uint8_t si = pdec->mstart; si > 0; --si)
		{
			mc_sync_pattern[mc_sync_count] = pdec->pattern[pdec->message[si]];
			mc_sync_count++;
			if (mc_sync_count > 3) break;
		}
	}

#if DEBUGDETECT >= 1
		DBG_PRINTLN("  -- MC found -- ");
#endif
	return true;
```

Ralf9, 2017-12-09:
Wie sieht es mit Deiner Anpassung mit Somfy aus. Ich habe mal 3 Fälle aufgezeichnet.
Wird "Da" noch als "1" erkannt?
Bei Fall b wird "Db" wahrscheinlich als "1" erkannt, da dann aber "Dbf" ungültig ist, kann mstart um eins erhöht werden.
Wie sieht es mit Fall c aus, da müssten dann die ganzen Daten nach einem ungültigen MC-Puls durchsucht werden und dann mstart um eins erhöht werden und dann nochmals die Daten nach einem ungültigen MC-Puls durchsucht werden.

[image]

sidey79, 2017-12-09:
Hmm, Da würde ich auf den 1. Blick als 01 demodulieren.
Ob der SW Sync da etwas verändert weiss ich jetzt nicht. Das müssten wir einfach mal testen.

Fall B wird dbf gefunden und dann geht es zurück.

Fall c kann ich nicht sagen, da ich keinen Long Pulse sehe. Auf jeden Fall ist dca und dcb identisch

Im wesentlichen häng es davon ab, wo man den Beginn der Manchester Übertragung setzt. Das ist auch der Grund, wieso man eigentlich erst ab einem langen Puls anfangen kann zu demodulieren. Alles davor sind annahmen.
Woher soll man wissen, ob der syncpuls von einem Short oder long Puls mit gleichem Pegel verlängert wurde?

Ralf9, 2017-12-09:
> Fall B wird dbf gefunden und dann geht es zurück.

Bei Fall b und c dürfte es bei einem erkannten ungültigen MC-Puls nicht sofort zurück gehten.
Der Beginn der Manchester Übertragung müsste um eins erhöht werden und dann nochmals nach einem ungültigen MC-Puls gesucht werden. Wenn es dann passt, dann wurde der richtige Beginn gefunden

sidey79, 2017-12-09:
Wo ist denn bei B und C ein ungültiger MC Puls?

Ralf9, 2017-12-09:
wenn Db als 1 erkannt wird, ist dann Dbf ein gültiger MC-Puls.
Wenn beim Fall c Dca und Dcb als 1 erkannt werden, dann wird es beim nächsten Bitwechsel auch einen ungültiger MC Puls geben

Ralf9, 2017-12-09:
Wenn ich das richtig sehe darf nach einem SH kein LL folgen.

Ralf9, 2017-12-09:
Und hier habe ich mal das osv1 aufgezeichnet.
Die Prüfung "Bei einer erkannten 1 darf nach einem SH kein LL folgen" passt hier auch.
[image]

sidey79, 2017-12-09:
Ich weiss nicht genau was Du meinst.

Die Bits werden immer durch zwei Pegelstände moduliert.

Entweder wird ein low und dann einem high Puls generiert oder halt umgedreht erst high und dann low.

Durch das zusammensetzen einer 1 und einer 0 entsteht dann das, was wir als langen Puls bezeichnen. Auf der Modulationsseite gibt es aber nur Pulse im Takt.
Deine Annahme, dass nach einem SH kein LL folgen dürfe, kann ich nicht nachvollziehen. 
Das Passiert meiner Meinung nach, wenn man eine 0 und dann eine 1 Moduliert .

Richtig ist, dass ein kurze Pulse immer paarweise auftreten müssen. Lange Pulse können in beliebiger Anzahl auftreten.

So ganz habe ich leider nicht verstanden, was Du mit dem erhöhen von mstart meinst.
Der Beginn des Signals wird relativ vom 1. langen Puls berechnet. Das ist der Punkt, an dem wir uns auf das Signal synchronisieren können.

Kompliziert ist, wenn der Pegelstand zwischen Ende eines Sync Pulses und Beginn der MC Nachricht identisch ist.
Alles in allem, muss man sich vor Augen führen, wie die MC Modulation beginnt. An das Sync Signal kann entweder der gleiche Pegelstand angehangen sein oder nicht.
Und nur eine Variante kann aufgehen, wenn man relativ vom Synchronisationspunkt (lange puls) ausgeht.

So habe ich es meiner Meinung nach implementiert.

Ralf9, 2017-12-09:
Mir ist nicht klar was Du mit dem 1. langen Puls meinst.

Beim Fall a beim osv1, ist der Beginn der low - high Wechsel nach dem low Sync (5000).
Beim Fall b passt dies mit dem low - high Wechsel nicht. Dba und Dbb passen noch, aber nach Dbb passt es nicht mehr. Nach dem SH vom Dbb darf kein LL folgen.
Wenn man nun den Beginn um einen Puls auf die fallende Flanke nach hinten schiebt, dann Passt es

sidey79, 2017-12-09:
Wenn man den Takt nicht kennt, kann man sich erst auf das Signal synchronisieren, wenn ein langer Pegelstand empfangen wurde.
Wenn Du du mir sagst, wie es gehen soll bin ich ganz ihr dafür.

Ausgehend von dem langen Puls kann man dann feststellen, wie man die davor empfangenen Signalstärke auswerten soll.
Um es noch Mal zu verdeutlichen, bevor ein langer Puls empfangen wurde, hat man nur identische Bits empfangen. Wenn man nicht weiss, wo die Moduliert  angefangen hat, weiss man nicht ob das 1en oder 0er sind.

Der 1. Lange Puls in deinem Beispiel osv1 Signal ist dbf

Ralf9, 2017-12-09:
> Der 1. Lange Puls in deinem Beispiel osv1 Signal ist dbf

Wenn Du on diesem Fall Dbf als Beginn nimmst, dann fehlen Dir aber die beiden "00" davor,

Ich würde bei einer Nachricht mit Sync erstmal bei dem low - high Wechsel nach dem low Sync anfangen. Bei dem ersten langen Puls weiß ich dann ob ich richtig angefangen habe.
Wenn ich feststelle, daß ich falsch angefangen habe, nehme ich den high - low wechsel

sidey79, 2017-12-09:
Hi Ralf,

Manchester kenn keinen Sync. Der ist nicht definiert. Die einzige Option ist, mithilfe eines langen Pulses sich auf das Signal zu synchronisieren.
Die Bits vor dem Puls auf den ich synchronisieren fehlen nicht, da ich den Start anschließend noch vom Sync Punkt aus suche. Also einfach eine Schleife rückwärts zählen lasse.

Ralf9, 2017-12-09:
Jetzt ist es mir klar wie Du es machst, Wenn Du es richtig programmiert hast müsste es eigentlich auch beim Somfy und osv1 zuverlässig funktionieren.
Beim Hideki passt es jetzt. Ich habe den Eindruck, daß wir damit die zusätzliche ID 12.1 nicht mehr benötigen. 
Seit wann hast Du das mit langen Puls zum syncronisieren eingebaut?

Evtl lässt sich beim finden vom Start noch was verbessern. Am Besten ich zeichne es auf wie ich es mir vorstelle.

sidey79, 2017-12-09:
Wenn das Wörtchen wenn nicht wäre :)

Ich habe das schon vor Ewigkeiten eingebaut, kann mich nicht erinnern wann :(

Gestern habe ich nur einen Fehler behoben.
Die Preamble Erkennung war in einem Define fürs Debugging gefangen.

Ich vermute dass die Polarität nun wieder so wie früher ist, wenn eine Preamble erkannt wurde.

Ralf9, 2017-12-09:
Ich habe die Erkennung von 1.Bit mal aufgezeichnet.
[image]

Ralf9, 2017-12-09:
Demnach können vor dem ersten langen low Puls nur Nullen und vor dem ersten langen high Puls nur Einsen.

sidey79, 2017-12-09:
Ja, ich glaube so ist es. Ich muss da selbst auch erst immer mehrmals umdenken an welcher Stelle der Pegelzustand das Bit bestimmt. :)

Ralf9, 2017-12-09:
Ich habe mir mal die Routine angeschaut, die den Start von Syncpunkt rückwärts sucht, wenn ich es richtig überblicke, werden damit bei meiner Zeichnung nur Fall B und D berücksichtigt.

```
		if (mc_sync == false && (isLong(pdec->message[i])))
		{
			if (i>0) // Todo: Prüfen ob das 1. Bit damit korrekt ermittelt wird
				bit = pdec->message[i] == longhigh ? 0: 1; 
			mc_sync = true;
			pdec->mstart = i;  // Save sync position for later
			//if (i > 2) i=i-2;

			// Check if ther are min three more short pulses bevore
			if (i > 2 && isShort(pdec->message[i -3]) && isShort(pdec->message[i - 2]) &&  isShort(pdec->message[i - 1]) )
			{
				i--;  // Todo: Alle bereits erkannten Pulse zurücksüpringen
				prelongdecoding = true;
				bit = bit ^ 1; // need to flip the bit 
			}
			while (i>1 && isShort(pdec->message[i - 1]) && isShort(pdec->message[i - 2]))
			{
				i = i - 2;
			}
		}
const uint8_t mpi = pdec->message[i]; // Store pattern for further processing
```

Ralf9, 2017-12-09:
Möchstest Du mit der preamble Fälle ähnlich Fall a abdecken, wenn der Syncpunkt high ist.
Dann müsste, wenn der Syncpunkt low ist, keine peamble behandlung notwendig sein.

sidey79, 2017-12-10:
Ja. Fall Aber war der Anlass für die Preamble Erkennung.
Ich weiss nur nicht mehr, wer das implementiert hat. Müsste ich jetzt in den commitet suchen.
Ich habe es nur angepasst, damit es auch mit den anderen Änderungen funktioniert.

Ralf9, 2017-12-10:
Ich finde so wie es momentan mit der Preamble Erkennung gelöst ist, ist es recht schwer zu durchschauen und nach Fehlern zu suchen.
Es ist wahrscheinlich am Besten die Preamble Erkennung rauszunehmen und es dann sauber zu progammieren, daß alle Fälle berücksichtigt werden.
Wenn ich mir den o.g. Code anschaue, ist es mir klar, daß damit osv1 und Somfy nicht sauber funktioniert.

sidey79, 2017-12-10:
Mag sein, wenn Du eine andere Idee hast, dann lass es mich wissen.

Die Erkennung prüft halt aktuell, ob der letzte Preamble Puls einem vielfachen der Taktrate entspricht :)

Ralf9, 2017-12-11:
Mir ist nicht klar warum die Preamble Erkennung einen einfluss darauf hat ob die Nachricht invertiert ist.
Wenn ich das richtig kapiere wird doch durch den ersten langen Puls die Polarität bestimmt.
Wenn der erste lange Puls LOW ist, dann sind alle Bits davor 0 und es ist keine Preamble Erkennung notwendig.
Wenn der erste lange Puls HIGH ist, dann sind sind alle Bits davor 1.

sidey79, 2017-12-11:
Durch die Preamble ist der 1. Lange Puls eventuell in der Preamble enthalten.

Das müsste man über die Anzahl an Short Pulsen zum nächsten Long berechnen können.

Im ganzen dürfte es vier Situationen geben.
Die sollten wir in Form von Signalen definieren .

Dann können wir auch verifizieren wie es erkannt wird.
Mir fällt es schwer, mich da immer wieder rein zu denken.

Ralf9, 2017-12-11:
In der Protokollbescheibung von osv1 und somfy konnte ich kein long in der Preamble erkennen.
Und bei den anderen wie z.B. osv2 und 3 müsste es doch egal sein, wenn der 1. Lange Puls eventuell in der Preamble ist, da wir da auch die Preamble mit decodieren

sidey79, 2017-12-11:
Ja stimmt, einen Long kann es ja nicht geben, es kann nur die 1. Hälfte des Signals vom 1. Bit verbunden sein.

sidey79, 2017-12-12:
Ich hab jetzt auf einem Blatt Papier die verschiedenen Möglichkeiten aufgemalt.

Eigentlich spielt der letzte Puls vor einem validen Manchester Signal keine Rolle. Den müssen wir nach meinen Versuchen eigentlich nie auswerten. Wundert mich jetzt ein wenig, wieso das mal eingebaut wurde.

Ralf9, 2017-12-12:
Ich habe meine Zeichung um den Fall A2 ergänzt.
Es gibt 2 Fälle.
- wenn der lange Puls low ist (Fälle B - D), dann muß die Preamble nicht beachtet werden.

- wenn der lange Puls high ist (Fälle A1 u A2), dann ist es eine gültige 1, wenn der Puls "d" ein kurzer oder langer (SH oder LH) Puls ist. Die Länge von Puls "c" spielt dabei keine Rolle.

Ralf9, 2017-12-12:
Bei der Decodierung würde ich beim ersten langen Puls beginnen.
Zuerst vom langen Puls bis zum Start zurückgehen und dabei jeweils ein "ManchesterBits.addValue(bit);"

Dann ab dem langen Puls beginnen zu decodieren.
Eine äußere Schleife für die langen Pulse und eine innere Schleife für die kurzen.

Wenn nach einem langen low Puls ein langer high Puls kommt, dann weiter auf der Äußeren Schleife mit dem langen high Puls.
Wenn ein kurzer high Puls (SH) kommt, dann weiter in der inneren Schleife, wenn nach dem SH Puls
ein langer low (LL) Puls kommt, dann ist das Signal invallid und die decodierung wird beendet.
Bei einem kurzem low wird in der inneren Scheife weitergemacht.

sidey79, 2017-12-12:
Wenn ich es richtig verstanden habe, dann machen es die Schleifen fehleranfällig.

Vielleicht habe ich es aber auch falsch verstanden. Ich bin gerade am überlegen dass ich diese preamble Erkennung nicht brauche. das geht auch ohne.

sidey79, 2017-12-12:
Ich hab das mal vereinfacht und Fall A1 durchgespielt:

Alles wie gehabt, erst wird mal der erste lange Puls gesucht und in mstart gespeichert:
```
		if (mc_sync == false && (isLong(pdec->message[i])))
		{
			if (i>0) // Todo: Prüfen ob das 1. Bit damit korrekt ermittelt wird
				bit = pdec->message[i] == longhigh ? 0: 1; 
			mc_sync = true;
			pdec->mstart = i;  // Save sync position for later
```

Jetzt kann man problemlos für 2x short immer ein Bit speichern. Allerdings müssen wir einmal einen flip des bits machen, da der long ja gleichbedeutend mit einem flip ist.
```
			while (i>1 && isShort(pdec->message[i - 1]) && isShort(pdec->message[i - 2]))
			{
				if (i==pdec->mstart) bit = bit ^ 1; // need to flip the bit once
				i = i - 2;
				ManchesterBits.addValue(bit); // only short pulses, we can add them in reverse order
				
			}
```
Wir haben jetzt ein Bit im Bitpuffer (1), da wir wissen uns fehlt noch genau eines, können wir das einfach hinzufügen, solange wir rückwärts gegangen sind, muss das immer so sein:
```
			if (i < pdec->mstart) // We have moved back
			{
				i = pdec->mstart;
				ManchesterBits.addValue(bit); // add a bit for the first half of our long pulse
			}
```
Im Bitpuffer müsste dann 11 stehen..
Die Nächste Prüfung kann man problemlos auf das Setzen von mc_start_found reduzieren, mehr brauchen wir da nicht mehr.
```
		if (mc_start_found == false && mc_sync && (isShort(mpi) || isLong(mpi)))
		{
			pdec->mstart = i;
			mc_start_found = true;
```

An dem Hinzufügen der Bits muss man jetzt nichts ändern. Bei Langen Pulsen wird geflipt und bei zwei short pulsen wird ein bit hinzugefügt identisch zum letzten.
Beim 1. Durchlauf nachdem wir mc_sync gefunden haben, stehen wir mit der Variable I noch immer auf dem 1. langen Puls, also einmal das bit umdrehen:
```
			else if (isLong(mpi)) {
				bit = bit ^ (1);
                                hasbit=true;
			}
```
```
			if (mc_start_found) { // don't write if manchester processing was canceled
				if (hasbit) {
					ManchesterBits.addValue(bit);
```

Und schon haben wir 110 im Bitpuffer stehen.

sidey79, 2017-12-13:
@Ralf9 

Bin jetzt noch die anderen Varianten durchgegangen.
Abschnitt zwei musste ich noch etwas anpassen um auch die anderen Fälle abdecken zu können.

	```
		while (i>1 )
			{
				if (i == pdec->mstart) {
					bit = bit ^ 1; // need to flip the bit once
					if (abs(pdec->pattern[pdec->message[i - 1]]) > pdec->pattern[longhigh])
					{
						ManchesterBits.addValue(bit); // Add one bit to our buffer.
						break;
					}
				}
				if (isShort(pdec->message[i - 1]) && isShort(pdec->message[i - 2]))
				{
					i = i - 2;
				}
				ManchesterBits.addValue(bit); // only short pulses, we can add them in reverse order

			}
```

Mein Schreibtischtest kommt Bei A1, A2 und B auf das gleiche Ergebnis wie in deinem Schaubild.

Bei Somfy (C) kommt 001 und bei Somfy (D) 011 heraus.
Das ist meiner Meinung nach, auch korrekt so.

Man könnte das wohl noch etwas schicker lösen, aber das Ergebnis müsste schon mal so passen.

Ralf9, 2017-12-13:
Ich habe den Fall D geändert
[image]

Ralf9, 2017-12-13:
Wegen dem Fall Dneu muß die Suche nach dem ersten langen Puls geändert werden:
```
uint8_t pulseid = pdec->message[i];
if (mc_sync == false &&  (pulseid == longhigh || (pulseid == longlow && pdec->pattern[pdec->message[i-1] <= pdec->pattern[longhigh])
```

Ralf9, 2017-12-13:
Ich habe es ein wenig überarbeitet:
```
		while (i>1 )
		{
			if (i == pdec->mstart) {
				bit = bit ^ 1; // need to flip the bit once
				if (pulseid == longlow && pdec->pattern[pdec->message[i - 1]] > pdec->pattern[shortlow])
				{
					ManchesterBits.addValue(bit); // Add one bit to our buffer.
					break;
				}
				if (isShort(pdec->message[i - 1]))
				{
					i--;
				}
			}
			if (isShort(pdec->message[i -2]) && isShort(pdec->message[i - 1]))
			{
				i = i - 2;
			}
			else {
				if (pdec->message[i - 1] == shorthigh)  // Fall A1
				{
					ManchesterBits.addValue(bit); 
				}
				break;
			}
			ManchesterBits.addValue(bit); // only short pulses, we can add them in reverse order

		}
```

Ralf9, 2017-12-13:
[image]

sidey79, 2017-12-13:
Hi Ralf,

ein paar Fragen / Anmerkungen:
Wir sind hier aber ohnehin bei dem Fall, dass wir vorher geprüft haben ob i == pdec->mstart ist. pdec->mstart steht auf dem 1. Long. Ob high oder low spielt meiner Meinung nach keine Rolle. Ebenso die Prüfung auf shortlow, der Pegelstand ist egal und es muss größer einem long sein, sonst stimmt da doch etwas nicht.

`				if (pulseid == longhigh && pdec->pattern[pdec->message[i - 1]] > pdec->pattern[shortlow]) 
`

Ich frage mich, warum break wenn i<3...
```
				if (isShort(pdec->message[i - 1]))
				{
					if (i < 3 ) break;
					i--;
				}
```

Dieser Abschnitt kommt meiner Meinung nach bei A1 nicht zum tragen:
```
else {
				if (pdec->message[i - 1] == shorthigh)  // Fall A1
				{
					ManchesterBits.addValue(bit); 
				}
```

Bei Fall B glaube ich kommt mit deinen Anpassungen nur 01 in den Bitpuffer.

Mit dneu bin ich noch unsicher, den 1. langen puls zu ignorieren nur weil kein kurzer davor ist.

Ralf9, 2017-12-14:
dneu ist der Sonderfall Somfy, wo der letzte low sync shortlow ist. Den möchte ich der Einfachheit ausklammern. Bei dneu ist der Puls d kein richtiger long.

> if (pulseid == longhigh && pdec->pattern[pdec->message[i - 1]] > pdec->pattern[shortlow])

Sonst würde der Fall Dneu nicht passen und bei long low kommt immer davor ein short.

Die Abfrage if (i < 3 ) break brauche ich für den Fall nA1 und nA2 da muss mstart größer 2 sein

> if (pdec->message[i - 1] == shorthigh)  // Fall A1

Dies ist eine Abfrage auf Puls d

sidey79, 2017-12-14:
Okay, ich habe verstanden, Du hast die Bedingungen nach und nach an die Beispiele angepasst.
Ist eine valide Vorgehensweise.

Einen long Abhängig davon zu Behandeln ob er high oder low ist, ist meiner Meinung nach nicht notwendig. 
In den Beschreibungen die es zu Manchester gibt, habe ich aber nie solche Vorgehensweisen gesehen.

Ich verstehe durchaus, wenn man die Beispiele durchgeht, dass man dann zu Bedingungen kommt die genau ein Beispiel abdeckt. So bekommt man auch irgendwann alle Beispiele abgedeckt, allerdings kann dann jeder Sonderfall in einer eigenen Bedingung enden.  Bitte nicht falsch verstehen, ich persönlich würde noch versuchen diese Sonderbedingungen zu verallgemeinern.

sidey79, 2017-12-14:
Noch mal zu dneu:

Wenn ich 010 übertragen möchte, dann sieht das doch genau so aus, wie bei dneu. Aus meiner Sicht, würde man das als 010 dekodieren.

sidey79, 2017-12-14:
Ich hab noch mal ein paar Sachen umgestellt. Sollte so auch schneller sein und weniger code erzeugen.

Die Abfrage auf den longlow als Start ohne vorhergehenden shorthigh habe ich entfernt, da meiner Meinung nach dann das 1. Bit nicht erkannt wird, wenn es 0 ist und vor dem Manchester Signal der Pegel high war.
Die innere Schleife brauchen wir ja  nur, wenn vor unserem long ein short kommt und wir noch mindestens drei Pulse vorher finden

```
		if (mc_sync == false &&  isLong(pdec->message[i]) )
		{
			bit = pdec->message[i] == longhigh ? 0: 1; 
			mc_sync = true;
			pdec->mstart = i;  // Save sync position for later
	
			if (i >1) 
			{
				bit = bit ^ 1; // need to flip the bit once
				ManchesterBits.addValue(bit);
				if (isShort(pdec->message[i - 1]) && --i>2 )
				{
					while (i > 1 )
					{
						if (isShort(pdec->message[i -2]) && isShort(pdec->message[i - 1]))
						{
							i = i - 2;
						} else {
							if (pdec->message[i - 1] == shorthigh && pdec->pattern[pdec->message[i - 2]] < pdec->pattern[longlow])  // Fall nA1
							i=0; // leave the while loop
						}
						ManchesterBits.addValue(bit); 
					}
                                        i = pdec->mstart;
				} 
			}			

```

Ralf9, 2017-12-14:
Ich habe in den beiden Zeichnungen alle für mich denkbaren Fälle aufgezeichnet. Ich gehe davon aus, daß ein soft sync immer mit einem low Puls endet.
Kennst Du Fälle die damit nicht abgedeckt sind. 

Ich habe es versucht soweit es geht zu verallgemeinern.
Deine Umstellung ist mir zu kompliziert, ich finde es so schwieriger nachzuvollziehen.
Da in der Regel nur recht wenige Pulse vor dem ersten long sind, dürfte die Geschwindigskeitoptimierung nicht so wichtig sein.

sidey79, 2017-12-14:
Also der Pegelstand vor Beginn des MC Signals kann entweder low oder high sein, das sind die Optionen die es zu berücksichtigen gibt.

Die Spezifischen Protokolle kann man zum Verifizieren nehmen, das ist schon gut.

Durch das Umstellen ist erkennbar, welche Codezeilen wann ausgeführt werden, da diese ja verschachtelt sind und nicht mehr so viele parallel laufen

sidey79, 2017-12-14:
Einziger Haken bei der Rückwärts Implementierung ist, dass wir die Debug Ausgabe nicht ordentlich hinbekommen

sidey79, 2017-12-20:
Habe noch zwei Sachen angepasst.

1. i>1 habe ich durch i>0 ersetzt. Passt also ab dem 2. Puls im Puffer.

```
			if (i >0 ) // Todo: Eventuell reicht auch i>0 ?
			{
				bit = bit ^ 1; // need to flip the bit once
				ManchesterBits.addValue(bit);
				
				if (isShort(pdec->message[i - 1]) && --i>2)
				{
					while (i > 1)
					{
						if (isShort(pdec->message[i - 2]) && isShort(pdec->message[i - 1]))
						{
							i = i - 2;
						} else {
							// Letzter Durchlauf
							if (pdec->message[i - 1] == shorthigh && pdec->pattern[pdec->message[i - 2]] < pdec->pattern[longlow]) 
							{
								i = 0; // leave the while loop after adding the value
							} else { 
								break;  // Add no more value
							}
						}
						ManchesterBits.addValue(bit);
					}
				}
				if (i == pdec->mstart) i++; // 1. long shoud not be processed twice if there was nothing valid before that pulse
				else i = pdec->mstart; // recover i to mstart
```

Am Ende habe ich i um eins erhöht, wenn vor dem 1. langen Puls nichts mehr gefunden werden konnte. Ansonsten wird der 1. Puls ja 2x verarbeitet. Da bin ich mir aber nicht ganz sicher ob das nicht irgendwo ein Problemchen verursacht.

Ralf9, 2017-12-20:
https://github.com/RFD-FHEM/SIGNALDuino/issues/75#issuecomment-353041459
> Woran ich im Moment noch überlege ist, ob man die Sendepause zwischen den Wiederholungen in irgend einer besonderen Art berücksichtigt.

Die Erkennung von Wiederholungen habe ich bei mir eingebaut. Ich mache es ähnlich wie bei den MS- und MU-Nachrichten
```
					if (p_valid || (!p_valid && (messageLen - mend) >= minMessageLen)) {  // wenn Nachrichtenende erkannt wurde, dann muss der Rest laenger als minMessageLen sein
						bufferMove(mend);
						mstart = 0;
						mcRepeat = true;
					}
```
Das "mcRepeat = true" verwende in doDecode.
Wenn bei doDecode nichts gefunden wird und ManchesterBits.valcount kleiner minbitlen ist, dann weiter suchen.
Wenn es keine Wiederholung ist und es wurde nach minMessageLen nichts gefunden -> weiter mit MU-Nachricht
```
		if (ManchesterBits.valcount < minbitlen)
		{
			if (pdec->mcRepeat == false && i > minMessageLen)	// es ist keine Wiederholung und es wurde nach minMessageLen nichts gefunden -> weiter mit MU-Nachricht
			{
				return false;
			}
			ManchesterBits.reset();		// weiter suchen
		}
		else 
		{
			pdec->mend = i;
			return true;
		}
```

Mir ist nicht klar warum Du für minbitlen als Default nur 17 gewählt hast.
Ich habe gesehen, das es ursprünglich mal 20 war
Ist Dir ein Mancester Protocol bekannt mit einer minbitlen kleiner 25?

sidey79, 2017-12-20:
@Ralf9 
Verstehe noch nicht ganz, was Du da gemacht hast. :(

Das Problem was ich erst mal gesehen habe ist, dass es "Preamble Pulse" gibt oder halt auch einfach eine Sendepause. Ich bin nicht mal sicher, ob beim OSV2 Protokoll die 32 Bits zu Beginn wirklich richtig moduliert wurden. Da gibt es ja nie einen kurzen Puls, das sieht eher so aus, als ob direkt mit einem langen angefangen wird. Ob jetzt in der Sendepause noch ein Bit 0 versteckt ist, weiss ich so halt auch nicht. 

Vielleicht kannst Du mir ja noch erklären, was dein Code bewirken soll.
Bezüglich der minBitLength, ja 17 gibt es, darum habe ich es reduziert. Es gibt auch noch kürzere Protokolle, daher habe ich es konfigurierbar gemacht.

Ralf9, 2017-12-20:
> Verstehe noch nicht ganz, was Du da gemacht hast. :(

Was ist nicht klar?
Durch die Abfrage bei `if (ManchesterBits.valcount < minbitlen)` habe ich die Hoffnung, daß damit bei manchen MU-Nachrichten, die fälschlicherweise als MC-Nachricht erkannt wurden, doch als MU-Nachricht ausgegeben werden, falls sie invaliden MC Code enthalten.

Es ist doch egal wie die Preamble anfängt, in der sub SIGNALduino_OSV2() wird ja nach dem sync gesucht. Die Preamble muss nur ausreichend lang sein
`if ($bitData =~ m/^.?(01){12,17}.?10011001/) `

sidey79, 2017-12-21:
Hi Ralf,  ich dachte deine Anpassungen sollen auf das Problem mit der Pause zwischen den Wiederholungen abzielen.

Wie die Preamble anfängt ist leider nicht egal, da wir 10011001 nicht finden, wenn wir am Anfang falsch demodulieren. Geanauer gesagt, geht die Preamble mit einer 1 zu ende und dann kommt ein sync byte, das 10011001 lautet.

> The RF pulse that ends at clock transition “-32” is not part of the preamble; it just exists to wake up the receiver and allow time for the AGC circuit to get adjusted. 

Damit ist der 1. High Puls gemeint, der am Anfang gesendet wird. Bei der 1. Übertragung finden wir den nicht. 

>  The first actual preamble bit is the “0” that occurs at clock transition “-31”. 

Preamble bei OSV2 fängt also mit 0 an.

> Now, take a look at the sync nibble following the preamble. In the next figure, the last bit of the last preamble bit pair occurs at clock transition “0”. The first bit of the first sync bit pair then occurs at transition “1”. Also, remember that the preamble sequence ends with a “01” pair. 

**Sync beginnt mit einer 1.**

Schauen wir noch mal die 1. Übertragung an:
`MC;LL=-1034;LH=922;SL=-547;SH=425;D=555533352D2CAACAAAACAB34B4B2AACB52AAAD0;C=487;L=153;R=0;`
Ich habe die 1. Bytes mal umgewandelt:
```
0101010101010101001100110011010100101101001011001010101011001010
```
Preamble endet mit:
10
Sync byte beginnt mit 01100110.

Lösung hier, vermutlich invertieren. Passt auch, wenn man die Preamble verlängert:

```
2017.12.21 12:07:16.810 4: THN132N_40_1 decoded Oregon: T: 26.6 BAT: ok
2017.12.21 12:07:16.806 5: OREGON: checksum1 = 57 berechnet: 57
2017.12.21 12:07:16.805 5: OREGON: sensor_id=ea4c BitsMsg=68 Bits=64
2017.12.21 12:07:16.803 5: OREGON: decoding delay=85 hex=44EA4C104068269003C
2017.12.21 12:07:16.797 5: dummyDuino: dispatch 44EA4C104068269003C
2017.12.21 12:07:16.794 5: dummyDuino Dispatch: 44EA4C104068269003C, -74 dB, dispatch
2017.12.21 12:07:16.767 5: dummyDuino Dispatch: 44EA4C104068269003C, test ungleich: disabled
2017.12.21 12:07:16.740 4: dummyDuino: OSV2 protocol converted to hex: (44EA4C104068269003C) with length (76) bits
2017.12.21 12:07:16.712 4: dummyDuino: OSV2 protocol detected: preamble_pos = 25
2017.12.21 12:07:16.686 5: dummyDuino: extracted data 10101010101010101010101011001100110010101101001011010011010101010011010101010101010100110101010011001011010010110100110101010101001101001010110101010101010100101111 (bin)
2017.12.21 12:07:16.659 4: dummyDuino: Found manchester Protocol id 10 clock 487 RSSI -74 -> OSV2o3
2017.12.21 12:07:16.631 4: dummyDuino/msg get raw: MC;LL=-1034;LH=922;SL=-547;SH=425;D=55555533352D2CAACAAAACAB34B4B2AACB52AAAD0;C=487;L=153;R=0;
```
Schauen wir uns also die Wiederholung an, da uns bei der 1. Übertragung ja doch einiges von der Preamble fehlt:

`MC;LL=-1045;LH=905;SL=-550;SH=425;D=AAAAAAAB332B4B4D54D5554D532D2D3554D2B5554A;C=487;L=167;R=0;
`

Sieht leider nicht so gut aus, wir haben die Preamble, aber nun ist das Invertieren schädlich:
```
2017.12.21 12:17:55.197 4: OREGON: ERROR: Unknown sensor_id=ab33 bits=72 message='48AB334100A199400E30'.
2017.12.21 12:17:55.196 5: OREGON: decoding delay=157 hex=48AB334100A199400E30
2017.12.21 12:17:55.189 5: dummyDuino: dispatch 48AB334100A199400E30
2017.12.21 12:17:55.187 5: dummyDuino Dispatch: 48AB334100A199400E30, -74 dB, dispatch
2017.12.21 12:17:55.160 5: dummyDuino Dispatch: 48AB334100A199400E30, test ungleich: disabled
2017.12.21 12:17:55.134 4: dummyDuino: OSV2 protocol converted to hex: (48AB334100A199400E30) with length (80) bits
2017.12.21 12:17:55.106 4: dummyDuino: OSV2 protocol detected: preamble_pos = 28
2017.12.21 12:17:55.080 5: dummyDuino: extracted data 010101010101010101010101010101001100110011010100101101001011001010101011001010101010101010110010101011001101001011010010110010101010101100101101010010101010101010110101 (bin)
2017.12.21 12:17:55.054 4: dummyDuino: Found manchester Protocol id 10 clock 487 RSSI -74 -> OSV2o3
2017.12.21 12:17:55.025 4: dummyDuino/msg get raw: MC;LL=-1045;LH=905;SL=-550;SH=425;D=AAAAAAAB332B4B4D54D5554D532D2D3554D2B5554A;C=487;L=167;R=0;
```

Wenn man es jetzt nicht invertiert passt es:

```
2017.12.21 12:17:57.278 4: THN132N_40_1 decoded Oregon: T: 26.6 BAT: ok
2017.12.21 12:17:57.275 5: OREGON: checksum1 = 57 berechnet: 57
2017.12.21 12:17:57.273 5: OREGON: sensor_id=ea4c BitsMsg=72 Bits=64
2017.12.21 12:17:57.271 5: OREGON: decoding delay=2 hex=48EA4C1040682690031C
2017.12.21 12:17:57.265 5: dummyDuino: dispatch 48EA4C1040682690031C
2017.12.21 12:17:57.263 5: dummyDuino Dispatch: 48EA4C1040682690031C, -74 dB, dispatch
2017.12.21 12:17:57.236 5: dummyDuino Dispatch: 48EA4C1040682690031C, test ungleich: disabled
2017.12.21 12:17:57.210 4: dummyDuino: OSV2 protocol converted to hex: (48EA4C1040682690031C) with length (80) bits
2017.12.21 12:17:57.182 4: dummyDuino: OSV2 protocol detected: preamble_pos = 31
2017.12.21 12:17:57.156 5: dummyDuino: extracted data 101010101010101010101010101010110011001100101011010010110100110101010100110101010101010101001101010100110010110100101101001101010101010011010010101101010101010101001010 (bin)
2017.12.21 12:17:57.129 4: dummyDuino: Found manchester Protocol id 10 clock 487 RSSI -74 -> OSV2o3
2017.12.21 12:17:57.101 4: dummyDuino/msg get raw: Mc;LL=-1045;LH=905;SL=-550;SH=425;D=AAAAAAAB332B4B4D54D5554D532D2D3554D2B5554A;C=487;L=167;R=0;
```

Das liegt meiner Meinung nach, an der langen low Pause. Wir könnten das jetzt versuchen im 00_SIGNALduino Modul zu kompensieren, allerdings haben wir dieses Problem ja auch bei anderen Protokollen. Da müsste uns noch was besseres einfallen..

sidey79, 2017-12-21:
Gut, habe das Signal mal mit und ohne die lange Pause am Anfang demoduliert.

In beiden Fällen kommt das identische Ergebnis heraus. Warum habe ich noch nicht verstanden. :(

Ralf9, 2017-12-21:
> ich dachte deine Anpassungen sollen auf das Problem mit der Pause zwischen den Wiederholungen abzielen

ja, mit den Anpassungen wird die Pause erkannt und die Wiederholungen ausgegeben.

Ich habe mit Deinen OSV2 Pulsen mal ein wenig getestet
https://github.com/RFD-FHEM/SIGNALDuino/issues/75#issuecomment-352909978
ich konnte damit bis jetzt Deine Probleme nicht nachvollziehen.

Ich habe dies gesendet
`SR;P2=-1034;P3=922;P4=-547;P5=425;P6=-9356;D=54323632323232323232323232345254345254345254323234523254323452325434523232323254345232323232323232325434523232325434525432345232543234523254345232323232325434523254323234523232323232323254323;`
und dann dieses Empfangen, was ok ist
`MC;LL=-1041;LH=908;SL=-559;SH=408;D=55555533352D2CAACAAAACAB34B4B2AACB52AAAD0;C=485;L=161;`

Ich habe dies gesendet
`SR;P2=-1034;P3=922;P4=-547;P5=425;P6=-9356;D=54323236323232323232323232345254345254345254323234523254323452325434523232323254345232323232323232325434523232325434525432345232543234523254345232323232325434523254323234523232323232323254323;`
und dann dies empfangen, was auch passt
```
mcResMpos=9   -> an der pos i=9 wurde ein ungültiger MC-Puls erkannt und dann weiter nach einem langen SyncPuls gesucht 
DMC;P0=632;P1=-432;P2=916;P3=-1040;P4=-9348;P5=-569;P6=407;D=701232324_232323232323232323256365256365256365232325632365232563236525632323232365256323232323232323236525632323236525636523256323652325632365256323232323236525632365232325632323232323232365232;e;
MC;LL=-1040;LH=916;SL=-569;SH=407;D=AAAAA666A5A5955955559566969655596A5555A;C=488;L=156;
```
Da "24" (916/-9348) ein MS-Sync ist musste ich den MS-Decoder deaktivieren.
Wenn ich dies einbaue wird, wenn MC Wiederholungen ausgegeben werden, die MS Decodierung uebersprungen:
```
		if (mcRepeat == false)
		{
			getClock();
			if (state == clockfound && MSenabled) getSync();
		}
		else	// wenn MC Wiederholungen ausgegeben werden, wird die MS Decodierung uebersprungen
		{
			state = detecting;
		}
```

sidey79, 2017-12-21:
Du hast mit SR gesendet, interessant ist halt was im Signalpuffer wirklich auf der Empfangsseite steht.
Senden und Empfangen ist meistens nicht identisch. Ich habe das ganze getestet in dem ich die beschrieben Signale in den Puffer geschoben habe.

Von einem echten V2 Sensor, wird ja ein langer high puls gesendet, bevor die eigentlich Preamble beginnt.

Ralf9, 2017-12-24:
Ich habe jetzt bei meiner Version  `#define SENDTODECODER 1`  zugefügt
damit wird in der send_raw Routine anstatt zu senden, die Pulse direkt dem Decoder übergeben.
Ich habe damit
`SR;P2=-1034;P3=922;P4=-547;P5=425;P6=-9356;D=3232323232323232323232323232323452543452543452543232345232543234523254345232323232543452323232323232323254345232323254345254323452325432345232543452323232323254345232543232345232323232323232543236323232323232323232323232323232345254345254345254323234523254323452325434523232323254345232323232323232325434523232325434525432345232543234523254345232323232325434523254323234523232323232323254323;`
direkt zum Decoder übergeben und folgendes erhalten:

```
DMC;P0=-32001;P1=922;P2=-1034;P3=-547;P4=425;P5=-9356;D=0_1212121212121212121212121212121342431342431342431212134212431213421243134212121212431342121212121212121243134212121243134243121342124312134212431342121212121243134212431212134212121212121212431215_121212121212121212121212121212134243134243134243121213421;O;
DMC;P0=-32001;P1=922;P2=-1034;P3=-547;P4=425;D=1_21212121212121212121212121212134243134243134243121213421243121342124313421212121243134212121212121212124313421212124313424312134212431213421243134212121212124313421243121213421212121212121243121;e;
```

```
2017.12.24 14:32:18.793 4 : sduinoE/msg READ: MC;LL=-1034;LH=922;SL=-547;SH=425;D=AAAAAAAA666A5A5955955559566969655596A5555A;C=487;L=168;s1;b1;w;
2017.12.24 14:32:18.793 4 : sduinoE: Found manchester Protocol id 10 clock 487 -> OSV2o3
2017.12.24 14:32:18.793 4 : sduinoE: OSV2 protocol detected: preamble_pos = 32
2017.12.24 14:32:18.793 4 : sduinoE: OSV2 protocol converted to hex: (44EA4C104068269003C) with length (76) bits

2017.12.24 14:32:18.893 4 : sduinoE/msg READ: MC;LL=-1034;LH=922;SL=-547;SH=425;D=55555554CCD4B4B2AB2AAAB2ACD2D2CAAB2D4AAAB4;C=487;L=167;s1;b1;
2017.12.24 14:32:18.893 4 : sduinoE: Found manchester Protocol id 10 clock 487 -> OSV2o3
2017.12.24 14:32:18.893 3 : sduinoE: OSV2 protocol detected: preamble_pos = 31
2017.12.24 14:32:18.893 4 : sduinoE: OSV2 protocol converted to hex: (44EA4C104068269003C) with length (76) bits
```
Ich kann Deine Probleme nicht nachvollziehen, das abtrennen und ausgeben der Wiederholung funktioniert wie gewünscht.

sidey79, 2018-02-07:
@Ralf9 
Bin dabei für jeden der Fälle eine Test zu schreiben.

Bei A2 klappt es leider nicht. Den A2 habe ich leicht verlängert, aber der Start ist das interessante:
`std::string dstr("MU;P0=-4200;P1=5700;P2=-6250;P3=3145;P4=-1260;P5=1673;P6=-2718;D=012345456;"); //10001
`

Dabei kommt folgendes heraus:

```
[07.02.2018 23:05:31 Informational] Sync:  Clock: NULL, Tol: 543, PattLen: 7 (7), Pulse: -2718, 1673, mStart: 0, MCD: 0, mtrunc: 0
[07.02.2018 23:05:31 Informational] Signal: 012345456.  [9]
[07.02.2018 23:05:31 Informational] Pattern:  P0: 1*[-4200] P1: 1*[5700] P2: 1*[-6250] P3: 1*[3145] P4: 2*[-1260] P5: 2*[1673] P6: 1*[-2718]
[07.02.2018 23:05:31 Informational] mlen:9:mstart: 0
[07.02.2018 23:05:31 Informational] P1S1S1L0:mpos=9:mstart=3:mend=9:vcnt=3:bfin:z:\benutzer\sven\documents\rfd-fhem\w32test\win32arduino-v1.3.39\testwin32arduino\tests.cpp(357): error: Expected equality of these values:
```

Das passiert, weil wir am Ende der while Schleife auf diese Bedingung stoßen:
`				if (i == pdec->mstart) i++; // 1. long shoud not be processed twice if there was nothing valid before that pulse
`

Der Fall A1 passt, den rest habe ich noch nicht erstellt.

sidey79, 2018-02-11:
@Ralf9 

Bei dem Test "B" klappts übrigens auch nicht:

` std::string dstr("MU;P0=-4200;P1=5700;P2=-6000;P3=1673;P4=-1260;P5=-2718;P6=3145;D=01234356;"); //001
`

Liegt meiner Meinung nach daran, dass immer zwei kurze Pulse notwendig sind. :(
```
[11.02.2018 17:24:13 Informational] Sync:  Clock: NULL, Tol: 629, PattLen: 3 (3), Pulse: 3145, -2718, mStart: 0, MCD: 0, mtrunc: 0
[11.02.2018 17:24:13 Informational] Signal: 012.  [3]
[11.02.2018 17:24:13 Informational] Pattern:  P0: 1*[1673] P1: 1*[-2718] P2: 1*[3145]
[11.02.2018 17:24:13 Informational] mlen:3:mstart: 0
[11.02.2018 17:24:13 Informational] :mpos=3:mstart=0:mend=3:vcnt=-1:bfin:z:\benutzer\sven\documents\rfd-fhem\w32test\win32arduino-v1.3.39\testwin32arduino\tests.cpp(379): error: Value of: result
[11.02.2018 17:24:13 Informational]   Actual: false
[11.02.2018 17:24:13 Informational] Expected: true
```

sidey79, 2018-02-11:
@Ralf9 
Mit commit 9f4a821   werden alle hier beschrieben Fälle nun wie gewollt demoduliert.

Den Fall, das die Wiederholte osv2 Nachricht z.B. mit AAA anfängt und die 1. Übertragung mit 555 habe ich nun endlich mal nachgestellt.
Das liegt doch tatsächlich an der Pause zwischen den Wiederholungen.

sidey79, 2018-02-12:
Man könnte die Pause zwischen den Wiederholungen löschen.
Das müsste Recht einfach gehen, wenn man prüft ob der Pulse < longlow ist und der Nächste Pulse Short oder long entspricht.

Wenn das zutrifft könnte man die Variable i vor dem buffermove einfach erhöhen.

Damit sollte sich das fehlerhafte Erkennen einer MS Nachricht erledigt haben.

Ralf9, 2018-02-12:
Ich habe mit meiner Version auch nochmals die Fälle A - C getestet. Es wird auch wie gewollt demoduliert.
s ist der sync und b ist der start.
```
Fall A1: 1100
SR;P0=-4200;P1=5700;P2=-6250;P3=3145;P4=-1260;P5=1673;P6=-2718;D=0125434545654363454545454545454545454545454545;
MC;LL=-2718;LH=3145;SL=-1260;SH=1673;D=C680000;C=1465;L=25;s5;b2;

Fall A2: 1000
SR;P0=-4200;P1=5700;P2=-6250;P3=3145;P4=-1260;P5=1673;P6=-2718;D=01234545654363454545454545454545454545454545;
MC;LL=-2718;LH=3145;SL=-1260;SH=1673;D=8D0000;C=1465;L=24;s3;b3;

Fall B: 0010 0011
SR;P0=-4200;P1=5700;P2=-5000;P3=3145;P4=-1260;P5=1673;P6=-2718;D=012545634545654363454545454545454545454545454545;
MC;LL=-2718;LH=3145;SL=-1260;SH=1673;D=2340000;C=1465;L=26;s6;b3;

Fall C: 0100 0010
SR;P0=-2416;P1=2416;P2=4550;P3=-604;P4=604;P5=-1208;P6=1208;D=010234563434345634343454343434343434343434;
MC;LL=-1208;LH=1208;SL=-604;SH=604;D=421FF8;C=603;L=21;s6;b5;
```

Ralf9, 2018-02-12:
> Den Fall, das die Wiederholte osv2 Nachricht z.B. mit AAA anfängt und die 1. Übertragung mit 555 habe ich nun endlich mal nachgestellt.

Es ist doch egal ob die Nachricht mit AAA oder 555 anfängt, dies ist doch das selbe, nur um 1Bit verschoben.

Ich kann nicht erkennen, daß die Pause zwischen den Wiederholungen stört.
Durch mcRepeat = true verhindere ich daß bei der MC-Wiederholung nach einem sync gesucht wird.
Die Wiederholung einer MC-Nachricht kann doch keine MS-Nachricht sein.

[2 more replies]

closed 2020-04-12 by sidey79