RFD-FHEM/SIGNALDuino #84: ME Funkklingel

maik-dmxc, 2018-01-21:
Hallo,

habe nun testweise einen SIGNALDuino zu laufen und will meine Türklingel einbinden.
Wo kann ich Hilfe zum auswerten oder verarbeiten bekommen? hier?

Empfang:

`2018.01.21 10:09:33 4 : RXArduino/msg READ: MU;P0=-432;P1=662;P2=-318;P3=-653;P4=337;P5=-5037;D=01212134542134213434213421212121345421342134342134213421212121345421342134342134212121213454213421343421342121212134542134213434213421212121345421342134342134212121213454213421343421342121212134542134213434213421212121345421342134342134212121213454213421;CP=4;O;
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: start pattern for MU Protocol id 16 -> Dooya shutter mismatches, aborting
2018.01.21 10:09:33 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: start pattern for MU Protocol id 27 -> remote27 mismatches, aborting
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: start pattern for MU Protocol id 28 -> IC Ledspot mismatches, aborting
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: Starting demodulation at Position 6
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.21 10:09:33 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: Starting demodulation at Position 2
2018.01.21 10:09:33 4 : RXArduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2018.01.21 10:09:33 5 : RXArduino: Starting demodulation at Position 5
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-650;LH=660;SL=-321;SH=340;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-656;LH=655;SL=-327;SH=334;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-672;LH=631;SL=-345;SH=322;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-657;LH=650;SL=-328;SH=335;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-655;LH=656;SL=-322;SH=338;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-655;LH=654;SL=-324;SH=338;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-653;LH=657;SL=-321;SH=340;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-653;LH=660;SL=-318;SH=341;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-660;LH=658;SL=-321;SH=338;D=DADAA8;C=329;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-656;LH=655;SL=-319;SH=341;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-655;LH=656;SL=-325;SH=339;D=DADAA8;C=329;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-654;LH=654;SL=-320;SH=338;D=DADAA8;C=327;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-649;LH=659;SL=-320;SH=339;D=DADAA;C=327;L=20;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-653;LH=662;SL=-321;SH=338;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-654;LH=657;SL=-319;SH=337;D=DADAA8;C=327;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-655;LH=652;SL=-325;SH=338;D=DADAA8;C=328;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-653;LH=656;SL=-321;SH=338;D=DADAA8;C=327;L=21;
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-655;LH=653;SL=-324;SH=337;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-653;LH=656;SL=-320;SH=340;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=653;SL=-326;SH=339;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=657;SL=-321;SH=342;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-655;LH=657;SL=-323;SH=340;D=DADAA8;C=329;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-664;LH=643;SL=-335;SH=328;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=662;SL=-319;SH=340;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-647;LH=664;SL=-317;SH=344;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-649;LH=661;SL=-316;SH=344;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-648;LH=658;SL=-318;SH=345;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-650;LH=659;SL=-318;SH=343;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=655;SL=-319;SH=342;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-653;LH=653;SL=-322;SH=341;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=650;SL=-327;SH=340;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-654;LH=654;SL=-321;SH=340;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-655;LH=661;SL=-318;SH=341;D=DADAA8;C=329;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=659;SL=-316;SH=343;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-650;LH=666;SL=-315;SH=341;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-650;LH=658;SL=-319;SH=341;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-649;LH=657;SL=-321;SH=341;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=654;SL=-323;SH=338;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-651;LH=655;SL=-320;SH=340;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-652;LH=651;SL=-327;SH=339;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-654;LH=659;SL=-317;SH=339;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-654;LH=649;SL=-328;SH=338;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-653;LH=654;SL=-321;SH=339;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-654;LH=660;SL=-321;SH=339;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-656;LH=653;SL=-320;SH=338;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-653;LH=662;SL=-318;SH=339;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-645;LH=659;SL=-317;SH=342;D=DADAA8;C=327;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-650;LH=662;SL=-317;SH=341;D=DADAA8;C=328;L=21;
2018.01.21 10:09:34 4 : RXArduino/msg READ: MC;LL=-667;LH=649;SL=-329;SH=341;D=DADAA8;C=330;L=21;`

was kann ich nun damit anfangen??

maik-dmxc, 2018-01-21:
Hi es ist definitiv die Klingel!

> Am besten ist es, wenn Du im Abstand von 10 Sekunden klingelst.
habe ich getan :)

ich hätte erwartet des ein unknown message kommt, damit ich ein doif daraus bauen kann.

mfg maik

maik-dmxc, 2018-01-21:
1.test
2018.01.21 09:04:14 4 : RXArduino/msg READ: MC;LL=-651;LH=658;SL=-317;SH=344;D=DADAA8;C=328;L=21;
2.test
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-655;LH=660;SL=-316;SH=333;D=DADAA8;C=327;L=21;
3.test
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-653;LH=660;SL=-318;SH=341;D=DADAA8;C=328;L=21;

maik-dmxc, 2018-01-21:
ab und zu verirrt sich ein falsches read dazwischen
2018.01.21 10:09:33 4 : RXArduino/msg READ: MC;LL=-649;LH=659;SL=-320;SH=339;D=DADAA;C=327;L=20;
2018.01.21 09:04:13 4 : RXArduino/msg READ: MC;LL=-669;LH=650;SL=-331;SH=320;D=DAD6B6AA;C=328;L=31;

maik-dmxc, 2018-01-21:
> Funken auf 433 MHz tut heutzutage ja doch sehr viel.

es gibt nur noch 2 weitere Geräte die hier funken.
1. Wetterstation_Aussenfühler
2. Rauchmelder

diese Daten werden aber nicht decodiert da sie vermutlich zu keinem bekannten Protokoll gehören.

maik-dmxc, 2018-01-21:
Die anderen Signale sehen so aus:
`2018.01.21 09:09:02 4 : RXArduino/msg READ: MU;P0=-548;P1=942;P2=-1988;P3=-1010;P4=1927;D=01213121312131212131312121313131313121212131313131312131242131213121312131213121312131213121312121313121213131313131212121313131313121312421312131213121312131213121312131213121213131212131313131312121213131313131213124213121312131213121312131213121312131;CP=1;O;
2018.01.21 09:12:02 4 : RXArduino/msg READ: MU;P0=-1984;P1=945;P2=-1011;P3=1922;D=01210121012101012121010101012101210101212121212121010103012101210121012101210121012101210121010121210101010121012101012121212121210101030121012101210121012101210121012101210101212101010101210121010121212121212101010301210121012101210121012101210121012101;CP=1;O;
2018.01.21 09:21:02 4 : RXArduino/msg READ: MU;P0=132;P1=-504;P2=944;P3=-1010;P4=-1983;P5=1924;P6=346;P7=-110;D=012324232423242324242323242423232323232424242323232323242324542324232423242324232423242324232423242423232424232323232324242423232323232423245423242324236767;CP=2;
2018.01.21 09:18:02 4 : RXArduino/msg READ: MU;P0=-1010;P1=940;P2=-1987;P3=1928;D=01012101212101012121212121010121210101010101012121232101210121012101210121012101210121012121010121212121210101212101010101010121212321012101210121012101210121012101210121210101212121212101012121010101010101212123210121012101210121012101210121012101212101;CP=1;O;

2018.01.21 09:12:02 4 : RXArduino/msg READ: MU;P0=-1982;P1=954;P2=-998;P3=1912;P4=292;P5=-140;P6=492;D=01212101010101210121010121212121212101010301210121012101210121012101210121012101012121010101012101245601012121212121210101;CP=1;
2018.01.21 09:15:02 4 : RXArduino/msg READ: MU;P0=946;P1=-1005;P2=-1983;P3=1936;P4=240;D=010201010202020201020102020101010101010202023201020102010201020102010201020102010202010102020202010201024;CP=0;
2018.01.21 09:18:02 4 : RXArduino/msg READ: MU;P0=-1009;P1=930;P2=-2002;P3=1916;D=01212121212101012121010101010101212123210121012101210121012101210121012101212101012121212121010121210101010101012121;CP=1;
2018.01.21 09:21:02 4 : RXArduino/msg READ: MU;P0=-324;P1=959;P2=-988;P3=-1967;P4=1944;P5=-26000;D=0121212121213121343121312131213121312131213121312131213131212131312121212121313131212121212131215;CP=1;
`

sidey79, 2018-01-21:
Ich habe da so einen Verdacht.

Kannst Du die Klingel noch einmal bei deaktivierten MCDecoder testen:

set disableMessagetype manchesterMC

maik-dmxc, 2018-01-21:
OK 

> config | MS=0;MU=1;MC=0

> 2018.01.21 16:45:32 4 : RXArduino/msg READ: MU;P0=-662;P1=651;P2=330;P3=-5033;P4=-329;D=01023141410232410241020241024141414102324102410202410232410241020241024141414102324102410202410241414141023241024102024102414141410232410241020241024141414102324102410202410241414141023241024102024102414141410232410241020241024141414102324102410202410241;CP=2;O;
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: start pattern for MU Protocol id 16 -> Dooya shutter mismatches, aborting
2018.01.21 16:45:32 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: start pattern for MU Protocol id 27 -> remote27 mismatches, aborting
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: start pattern for MU Protocol id 28 -> IC Ledspot mismatches, aborting
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: Starting demodulation at Position 1
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.21 16:45:32 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: Starting demodulation at Position 2
2018.01.21 16:45:32 4 : RXArduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2018.01.21 16:45:32 5 : RXArduino: Starting demodulation at Position 1
2018.01.21 16:45:32 4 : RXArduino/msg READ

maik-dmxc, 2018-01-21:
hier mal die 5 Empfangenen Signale

> 2018.01.21 16:45:32 4 : RXArduino/msg READ: [...]
2018.01.21 16:45:32 4 : RXArduino/msg READ: MU;P0=-322;P1=659;P2=-649;P3=340;P4=-5043;D=01012123430123012323012301010101234301230123230123010123012323012301010101234301230123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101;CP=3;O;
2018.01.21 16:45:32 4 : RXArduino/msg READ: MU;P0=-325;P1=656;P2=-656;P3=323;P4=-5031;D=01012301230123230123010101012343012301232301230101010123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123;CP=3;O;
2018.01.21 16:45:32 4 : RXArduino/msg READ: MU;P0=-5047;P1=334;P2=-656;P3=-323;P4=659;D=01012134212134213434343421013421342121342134343434210134210134213421213421343434342101342134212134213434343421013421342121342134343434210134213421213421343434342101342134212134213434343421013421342121342134343434210134213421213421343434342101342134212134;CP=1;O;
2018.01.21 16:45:32 4 : RXArduino/msg READ: MU;P0=-650;P1=658;P2=341;P3=-5033;P4=-322;D=01023141410232410241020241024141414102324102410202410232410241020241024141414102324102410202410241414141023241024102024102414141410232410241020241024141414102324102410202410241414141023241024102024102414141410232410241020241024141414102324102410202410241;CP=2;O;
2018.01.21 16:45:33 4 : RXArduino/msg READ: MU;P0=-320;P1=663;P2=-651;P3=334;P4=-5021;D=01012123430123012323012301010101234301230123230123010123012323012301010101234301230123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101;CP=3;O;
2018.01.21 16:45:33 4 : RXArduino/msg READ: MU;P0=-321;P1=658;P2=-654;P3=326;P4=-5026;D=01012301230123230123010101012343012301232301230101010123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123;CP=3;O;

Ralf9, 2018-01-21:
Dies ist recht kniffelig. Wenn ich das "#define minMessageLen" auf 24 erniedrige und die MU Nachricht an den Decoder sende

`MU;P0=-321;P1=658;P2=-654;P3=326;P4=-5026;D=01012301230123230123010101012343012301232301230101010123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123430123012323012301010101234301230123230123010101012343012301232301230101010123;CP=3;O;`

bekomme ich die MS-Nachrichten:

```
MS;P0=-325;P1=656;P2=-656;P3=323;P4=-5031;D=343012301232301230101010123230123010101012;CP=3;SP=4;
MS;P0=-325;P1=656;P2=-656;P3=323;P4=-5031;D=34301230123230123010101012;CP=3;SP=4;
MS;P0=-325;P1=656;P2=-656;P3=323;P4=-5031;D=34301230123230123010101012;CP=3;SP=4;
MS;P0=-325;P1=656;P2=-656;P3=323;P4=-5031;D=34301230123230123010101012;CP=3;SP=4;
MS;P0=-325;P1=656;P2=-656;P3=323;P4=-5031;D=34301230123230123010101012;CP=3;SP=4;
```
Dies passt so aber nicht ganz, es müsste so ausehen:
```
430123012323012301010101232301230101010123
43012301232301230101010123
43012301232301230101010123
43012301232301230101010123

43 01 23 01 23 23 01 23 01 01 01 01 23
```

Wenn ich an die MS-Nachricht hinten die 3 anhänge, bekomme ich damit

```
	"78"    => 			## Tuerklingel
        {
            name			=> 'doorring78',
			#comment		=> '',
			id          	=> '78',
			one		=> [-1,2],
			zero		=> [-2,1],
			sync		=> [1,-15,1],
			clockabs     	=> 325,
			format 		=> 'twostate',
			preamble	=> 'U78#',		# prepend to converted message	
			#clientmodule    => '',   	# not used now
			#modulematch     => '', 	# not used now
		},
```
```

2018.01.21 17:51:23.329 4 : sduinoD/msg get raw: MS;P0=-325;P1=656;P2=-656;P3=323;P4=-5031;D=3430123012323012301010101232301230101010123;CP=3;SP=4;
2018.01.21 17:51:23.330 4 : sduinoD: Matched MS Protocol id 78 -> doorring78, bitLen=20
2018.01.21 17:51:23.330 4 : sduinoD: Decoded MS Protocol id 78 dmsg U78#A5E5E length 20
```

sidey79, 2018-01-21:
Es ist aber kein MS Signal, so wie MS heute definiert ist

Ralf9, 2018-01-21:
Da der erste Teil der Nachricht größer als die minMessageLen 40 ist, müsste die Türklingel eigentlich als MS-Nachricht empfangen werden.
Dazu muß mit
set enable Messagetype syncedMS
der MS-decoder aktiviert werden.

sidey79, 2018-01-22:
@ralf
Der Decoder ist doch schon im Default aktiv und muss nicht explizit aktiviert werden.

Ich überlege gerade zwei Dinge.
1. Wie kriege ich dem MC Decoder angewöhnt diese Signale zu demodulieren.

2. Ist das Prüfen auf minMessageLen ein gutes Kriterium oder eher eine Krücke.
Je kürzer das Signal je höher ja die Anzahl an Wiederholungen die in den Puffer passen. So erkennt unser Auge, dass es sich um ein Signal handelt.
Vielleicht kann ich in Zukunft das Muster einer Übertragung abspeichern und dann vergleichen.

Bis da hin, ist wird es von allen FW Versionen wohl als MU ausgewertet, was auch nicht falsch ist.

Ralf9, 2018-01-22:
> config | MS=0;MU=1;MC=0

In diesem Fall ist der MS Decoder aber deaktiviert!

sidey79, 2018-01-22:
Hmm, beim 1. Post kennen wir die Config nicht. Ich gehe von Default aus.

maik-dmxc, 2018-01-22:
also

wenn alles an = config | MS=1;MU=1;MC=1
kommt nur das

> 2018.01.22 15:46:18 4 : RXArduino/msg READ: MC;LL=-648;LH=658;SL=-325;SH=338;D=6D6D54;C=328;L=22;

was soll an oder aus zum testen?

Ralf9, 2018-01-22:
@maik-dmxc 
falls es für Dich nicht zuviel Aufwand ist, könntest Du bitte die Funkklingel auch mal mit meiner  V 3.3.2-dev testen (MS=1;MU=1;MC=1)?
https://forum.fhem.de/index.php/topic,82379.0.html

und auch mit der 3.3.1 RC2 (MS=1;MU=1;MC=1)
https://forum.fhem.de/index.php/topic,58396.msg740610.html#msg740610

maik-dmxc, 2018-01-24:
OK 

mit 

> version | V 3.3.2-dev SIGNALduino - compiled at Jan  7 2018 19:27:09

> config | MS=1;MU=1;MC=1;Mred=1;Mdebug=1_MScnt=3;MuSplitThresh=7000;MdebFifoLimit=80

kommt nix mehr durch

> 2018.01.24 16:00:05 4 : RXArduino/msg READ: Mu;���;���;��;��;�Ȃ;���;���;d#A!Q!Q!Q!QQ!Q!!!Q!Q!!!!!!!!!!QVQ!Q!Q!Q!Q!Q!Q!Q!Q!QQ!Q!!!Q!Q!!!!!!!!!!QVQ!Q!Q!Q!Q!Q!Q!Q!Q!QQ!Q!!!Q!Q!!!!!!!!!!QVQ!Q!Q!Q!Q!Q!Q!Q;C1;O;
2018.01.24 16:00:05 4 : RXArduino/msg READ: Mu;��;���;���;���;��;���;���;d!!!!!!#E&!!!!!!!!!!!!!!!;C1;e;

Ralf9, 2018-01-24:
Mred=1; bedeutend, dass die messagekommpression eingeschaltet ist.
Mit get sduino RAW CDR kannst Du sie ausschalten.
Hast du noch eine etwas ältere 00_Signalduino.pm?

Ralf9, 2018-01-24:
Mu bedeutet, dass es eine komprimierte MU Nachricht ist.
Bei der aktuellen 00_Signalduino.pm wird dies erkannt und unkomprimiert ausgegeben

maik-dmxc, 2018-01-24:
Weiter

mit

> V 3.3.1-RC3 SIGNALduino - compiled at Jan 22 2018 22:09:46

> config: MS=1;MU=1;MC=1;Mred=1
garnix

> config: MS=0;MU=0;MC=1;Mred=1

> RXArduino/msg READ: DMc;P0=658;P2=-323;P3=-655;P4=334;D=01020203454203420343420342020202034542034203434203420202020345420342034342034202020203454203420343420342020202034542034203434203420202020345420342034342034202020203454203420343420342020202034542034203434203420202020345420342034342034202020203454203420343;O;

maik-dmxc, 2018-01-24:
> Hast du noch eine etwas ältere 00_Signalduino.pm?

keine Ahnung??

Ralf9, 2018-01-24:
Wenn du mit "get sduino RAW CDR" die messagekommpression ausschaltet ist dies egal

maik-dmxc, 2018-01-24:
Also das Ausschalten von MU bringt nix.

es kommen keine Antworten im log wenn ich die Klingel drücke!
die geloggten Einträge stammen warscheinlich von den Rauchmeldern.

erst wenn ich wieder die originale FW V 3.3.0 SIGNALduino - compiled at Sep 18 2016 00:28:28
aufspiele kommt die Klingel durch.

mit beiden anderen FW's 
V 3.3.1-RC3 SIGNALduino - compiled at Jan 22 2018 22:09:46
V 3.3.2-dev SIGNALduino - compiled at Jan 7 2018 19:27:09

zeigt mein nano keine Reaktion auf die Klingen, nur sporadisch auf die Rauchmelder/FunkTempSensor

mfg Maik

Ralf9, 2018-01-24:
Ich meinte nicht das Ausschalten von MU sondern nur die messagekommpression.
Es sieht dann so aus:
config: MS=1;MU=1;MC=1;Mred=0;

Ralf9, 2018-01-24:
Hast Du nähere Infos zu der Funkklingel?
Hat der Türsender DIP-Schaler?
Muss der Türsender an dem Empfänger angelernt werden?

maik-dmxc, 2018-01-24:
> Ich meinte nicht das Ausschalten von MU sondern nur die messagekommpression.
Es sieht dann so aus:
config: MS=1;MU=1;MC=1;Mred=0;

das meinte ich auch :)

Ralf9, 2018-01-24:
Es müsste eigentlich ab und zu als MS Nachricht erkannt werden.
Wie sieht es mit der folgenden konfig aus?
config: MS=0;MU=1;MC=0;Mred=0;

maik-dmxc, 2018-01-24:
also die Klingel ist eine drahtgebundene m-e Vistus AD +  mit zusätzlichem Funk (433Mhz) Sende Modul m-e VTX BELL.
als Empfänger dienen je eine, m-e BELL 200RX & m-e BELL 220.2RX
[m-e BELL 200RX.pdf](https://github.com/RFD-FHEM/SIGNALDuino/files/1660785/m-e.BELL.200RX.pdf)
[m-e BELL 220.2 RX.pdf](https://github.com/RFD-FHEM/SIGNALDuino/files/1660786/m-e.BELL.220.2.RX.pdf)
[m-e VISTUS AD-4010.pdf](https://github.com/RFD-FHEM/SIGNALDuino/files/1660787/m-e.VISTUS.AD-4010.pdf)
[m-e VTX-BELL.pdf](https://github.com/RFD-FHEM/SIGNALDuino/files/1660788/m-e.VTX-BELL.pdf)

maik-dmxc, 2018-01-24:
es wäre hilfreich wenn Ihr in fhem ein event auslösen könntet, wenn etwas empfangen wird und ggf. die raw daten in fhem als reading übergeben könntet.

dies würde mir zeigen das der nano überhaupt etwas aufgeschnappt hat, obs nun demoduliert werden kann oder nicht.

man kann das log im event monitor leider nicht filtern lassen.
und bei mir kommen massig Daten an im log.
dadurch könnte ich auch etwas übersehen.

maik-dmxc, 2018-01-24:
> 

Es müsste eigentlich ab und zu als MS Nachricht erkannt werden. Wie sieht es mit der folgenden konfig aus? config: MS=0;MU=1;MC=0;Mred=0;
--

welche FW meinst du??

Ralf9, 2018-01-24:
V 3.3.2-dev SIGNALduino

maik-dmxc, 2018-01-24:
so

FW 

> V 3.3.2-dev SIGNALduino - compiled at Jan 7 2018 19:27:09

config

> MS=0;MU=1;MC=0;Mred=0;Mdebug=1_MScnt=3;MuSplitThresh=7000;MdebFifoLimit=80

Signal

> 2018.01.24 18:35:38 4 : RXArduino/msg READ: MU;P0=-653;P1=325;P2=-5036;P6=-322;P7=658;D=67010167016767676701216701670101670167676767012167016701016701676767670121670167010167016767676701216701670101670167676767012167016701016701676767670121670167010167016767676701216701670101670167676767012167016701016701676767670121670167010167016767676701;CP=1;O;
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 16 -> Dooya shutter mismatches, aborting
2018.01.24 18:35:38 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 27 -> remote27 mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 28 -> IC Ledspot mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: Starting demodulation at Position 2
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.24 18:35:38 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: Starting demodulation at Position 2
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: Starting demodulation at Position 1
2018.01.24 18:35:38 4 : RXArduino/msg READ: MU;P0=-5024;P1=331;P2=-324;P3=659;P4=-656;D=01234123414123412323232341012341234141234123232323410123412341412341232323234101234123414123412323232341012341234141234123232323410123412341412341232323234101234123414123412323232341012341234141234123232323410123412341412341232323234101234123414123412323;CP=1;O;
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 16 -> Dooya shutter matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 16 -> Dooya shutter mismatches, aborting
2018.01.24 18:35:38 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 27 -> remote27 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 27 -> remote27 mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 28 -> IC Ledspot matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 28 -> IC Ledspot mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 31 -> pollin isotronic matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: Starting demodulation at Position 4
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.24 18:35:38 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: Starting demodulation at Position 2
2018.01.24 18:35:38 4 : RXArduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2018.01.24 18:35:38 5 : RXArduino: Starting demodulation at Position 3
[... cut ...]
2018.01.24 18:35:39 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.24 18:35:39 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.24 18:35:39 4 : RXArduino: Fingerprint for MU Protocol id 40 -> romotec matches, trying to demodulate
2018.01.24 18:35:39 5 : RXArduino: Starting demodulation at Position 2
2018.01.24 18:35:39 4 : RXArduino: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2018.01.24 18:35:39 5 : RXArduino: Starting demodulation at Position 3

Ralf9, 2018-01-24:
Bei der Version im normalen fhem update (SVN)
` $Id: 00_SIGNALduino.pm 15471 2017-11-21 23:34:50Z Sidey $`
gibt es das Attribut rawmsgEvent
"When set to "1" received raw messages triggers events"

maik-dmxc, 2018-01-24:
> Bei der Version im normalen fhem update (SVN)
$Id: 00_SIGNALduino.pm 15471 2017-11-21 23:34:50Z Sidey $
gibt es das Attribut rawmsgEvent
"When set to "1" received raw messages triggers events"

OK wie kann ich rausfinden ob ich diese *.pm habe??

Ralf9, 2018-01-24:
In deinem FHEM Verzeichnis gibt es eine 00_SIGNALduino.pm, die Version steht in der ersten Zeile.

Du bekommst die Version auch angezeigt, wenn Du "version" oben ins Kommandoeingabefeld eingibst.

maik-dmxc, 2018-01-24:
> 00_SIGNALduino.pm 13215 2017-01-23 20:09:44Z Sidey

OK :) wie updaten? einfach laden und austauschen?

Ralf9, 2018-01-24:
Einfach ein normales fhem update machen

maik-dmxc, 2018-01-24:
mach ich grade :) Danke 1 Jahr alt 

es kommen jetzt auch messages anderer Geräte (Rauchmelder/FunkThermometer) rein

> 2018.01.24 19:04:16 4 : RXArduino/msg READ: MU;P0=-302;P1=264;P2=-449;P4=467;P6=184;P7=-124;D=0121012401040404040404062126240474010101;CP=1;e;
2018.01.24 19:04:16 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 20 -> livolo matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: Starting demodulation at Position 3
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.24 19:04:16 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.24 19:04:16 4 : RXArduino/msg READ: MU;P0=1241;P1=-206;P2=474;P3=-314;P4=319;P6=-152;P7=-115;D=1212321434343214321232121262627210626210126270627212343;CP=2;
2018.01.24 19:04:16 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 20 -> livolo matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: Starting demodulation at Position 4
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.24 19:04:16 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.24 19:04:16 4 : RXArduino/msg READ: MU;P0=-344;P1=243;P2=-521;P3=126;P4=182;P6=444;P7=-260;D=010121232424212421242121232323212323210121010671;CP=1;
2018.01.24 19:04:16 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 20 -> livolo matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: Starting demodulation at Position 7
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 24 -> visivon remote matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 24 -> visivon remote mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.24 19:04:16 5 : RXArduino: applying filterfunc SIGNALduino_compPattern
2018.01.24 19:04:16 4 : RXArduino/msg READ: MU;P0=-295;P1=440;P2=-168;P3=243;P5=-449;P6=-124;D=01230103010353030121030301035303535353036;CP=3;e;
2018.01.24 19:04:16 5 : RXArduino: applying filterfunc SIGNALduino_filterSign
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 20 -> livolo matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: Starting demodulation at Position 6
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 29 -> HT12e remote matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 29 -> HT12e remote mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 30 -> unitec47031 matches, trying to demodulate
2018.01.24 19:04:16 5 : RXArduino: start pattern for MU Protocol id 30 -> unitec47031 mismatches, aborting
2018.01.24 19:04:16 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
[... cut ...]
2018.01.24 19:04:17 4 : RXArduino: Fingerprint for MU Protocol id 34 -> unknown34 matches, trying to demodulate
2018.01.24 19:04:17 5 : RXArduino: start pattern for MU Protocol id 34 -> unknown34 mismatches, aborting
2018.01.24 19:04:17 4 : RXArduino: Fingerprint for MU Protocol id 37 -> weather37 matches, trying to demodulate
2018.01.24 19:04:17 5 : RXArduino: start pattern for MU Protocol id 37 -> weather37 mismatches, aborting
2018.01.24 19:04:17 5 : RXArduino: applying filterfunc SIGNALduino_compPattern

was was ist kann ich noch nicht sagen.

Ralf9, 2018-01-24:
zum Schauen was jetzt alles empfangen wird, kannst Du jetzt wieder alles einschalten.

Bitte poste hier nur die Sachen die die Funkklingel betreffen.
Für weitere unbekannte Sensoren kannst ggf ein neues Issue aufmachen.

Die MU Nachrichten von der Funkklingel sind erst mal ausreichend, daß bei config: MS=1 nichts ausgegeben wird kann ich bei mir nachvollziehen.
Ich muss es mir erst mal in Ruhe anschauen, kann aber ein paar Tage dauern.

elektron-bbs, 2018-01-24:
Wenn er dann noch MC wieder einschaltet, könnte sich das Thema erledigt haben. Das Protokoll habt ihr wohl fast genau vor einem Jahr eingebunden:
[Funk-Gong Sender in fhem einbinden](https://forum.fhem.de/index.php/topic,64251.0.html)

maik-dmxc, 2018-01-25:
so

MC wieder angemacht, mit beiden FW's 331 & 332 .
wird nicht demoduliert.

habe also zum Test eine normale FunkSteckdose getestet, wird jedesmal ein anderes unknown erkannt.

Ich werde den Funkempfänger tauschen gegen einen CC1101, ist bereits bestellt und warte nur noch auf Lieferung.

mfg maik

sidey79, 2018-01-25:
Am Empfänger liegt es eher nicht.

Rxb6 oder rxb8 funktionieren schon ganz gut.

Jeder hat halt vor oder Nachteile.

sidey79, 2018-01-25:
Am Empfänger liegt es eher nicht.

Rxb6 oder rxb8 funktionieren schon ganz gut.

Jeder hat halt vor- und Nachteile.

Ralf9, 2018-01-25:
Der Rxb6 empfängt sehr gut.
Damit Funkklingel auch als MU-Nachrichten empfangen wird wenn alles an ist, sind noch ein paar Anpassungen in der Firmware notwendig. Ich habe schon einen Verdacht an was es liegt.
Wenn MS=1 ist müssten Intertechno kompatible Fernbedienungen problemlos erkannt werden.

Ralf9, 2018-01-27:
Ich habe in der Firmware was angepasst 
https://github.com/Ralf9/SIGNALDuino/commit/b3d39bbf1c2c8ad5069646528736e15151a399ee

Damit funkitioniert es auch, wenn alles an ist.
Wenn bei einer erkannten MS-Nachricht die Länge einer Nachricht kürzer wie die minMessageLen ist, dann wird sie als MU-Nachricht ausgegeben.

Hier ist die Firmware für den nano ohne cc1101
```
set sduino flash https://raw.githubusercontent.com/Ralf9/SIGNALDuino/dev-r332_cc1101/firmware/SIGNALduino_nano_332r.hex
```
Version:
`V 3.3.2-dev SIGNALduino - compiled at Jan 27 2018 01:05:21`

Es ist noch eine neue Protokolldefinition notwendig, diese reiche ich später nach.

sidey79, 2018-01-27:
@Ralf9 
Hast Du mal andere Signale  getestet, ob diese von der Anpassung beeinträchtigt werden?
Bin gerade dabei, die Bibliothek in einem win32 Projekt ans Laufen zu bekommen, damit ich endlich gut debuggen kann.

Ralf9, 2018-01-28:
Ich habe es nur mit Signalen mit den folgenden IDs getestet: 0,3,7,9,12 und 61

maik-dmxc, 2018-01-28:
Hi,

habe nun deine neue FW drauf & einen Rxb6 dran zu hängen.

> version | V 3.3.2-dev SIGNALduino - compiled at Jan 27 2018 01:05:21

> config | MS=1;MU=1;MC=1;Mred=0;Mdebug=1_MScnt=3;MuSplitThresh=7000;MdebFifoLimit=80

> 2018-01-28 10:57:51 SIGNALduino RXArduino RAWMSG MU;P0=656;P1=-656;P2=335;P3=-326;P4=-5024;D=01230121230123030303012423012301212301230303030124230123012123012303030301242301230121230123030303012423012301212301230303030124230123012123012303030301242301230121230123030303012423012301212301230303030124230123012123012303030301242301230121230123030303;CP=2;O;

mfg maik

Ralf9, 2018-01-28:
Evtl musst Du bei Dir sduino durch den namen von Deinem sduino (z.B. RXArduino) ersetzen

maik-dmxc, 2018-01-28:
> Evtl musst Du bei Dir sduino durch den namen von Deinem sduino (z.B. RXArduino) ersetzen

Ich weiß, mein Befehl lautete `set RXArduino flash https://raw.githubusercontent.com/Ralf9/SIGNALDuino/dev-r332_cc1101/firmware/SIGNALduino_nano_332r.hex`

Ralf9, 2018-01-28:
Hier ist die neue Protokolldefinition mit der ID 79
```
	"79" => ##  MU;P0=656;P1=-656;P2=335;P3=-326;P4=-5024;D=01230121230123030303012423012301212301230303030124230123012123012303030301242301230121230123030303012423012301212301230303030124230123012123012303030301242301230121230123030303012423012301212301230303030124230123012123012303030301242301230121230123030303;CP=2;O;
			# https://github.com/RFD-FHEM/SIGNALDuino/issues/84
		{
			name		=> 'VTX-BELL_Funkklingel',
			#comment	=> '',
			id		=> '79',
			zero		=> [-2,1], 	#
			one		=> [-1,2],   	# 
			start  		=> [-15,1],	#
			clockabs     	=> 330,
			format 		=> 'twostate',	    # 
			preamble	=> 'U79#',	# prepend to converted message	
			#clientmodule    => '',   	# not used now
			#modulematch     => '^TX......', # not used now
			length_min      => '12',
			#length_max      => '44',
		},

```
https://github.com/Ralf9/RFFHEM/commit/c18b1e5bd6bc1d9f12f877af9d6a7fd9559da7cc

Du kannst auch übergangsweise, bis sidey es bei sich eingebaut hat, auch meine 00_SIGNALduino.pm verwenden:
https://raw.githubusercontent.com/Ralf9/RFFHEM/master/FHEM/00_SIGNALduino.pm

Ralf9, 2018-01-28:
@maik-dmxc 
Von wann ist Deine 00_SIGNALduino.pm (steht in der 2.Zeile der 00_SIGNALduino.pm)?

maik-dmxc, 2018-01-28:
hab malk deine draufgespielt.
meine war vom fhem update vom 24.01.2018

werde es nachher mal testen.

mfg maik

maik-dmxc, 2018-01-28:
ups, habe ausversehen ge closed ?

hier das ergebniss mit deinem *.pm

`2018.01.28 19:14:52 4 : RXArduino: Fingerprint for MU Protocol id 79 -> VTX-BELL_Funkklingel matches, trying to demodulate
2018.01.28 19:14:52 5 : RXArduino: Starting demodulation at Position 23
2018.01.28 19:14:52 5 : RXArduino: dispatching bits: 1 0 1 0 0 1 0 1 1 1 1 0
2018.01.28 19:14:52 4 : RXArduino: decoded matched MU Protocol id 79 dmsg U79#A5E length 12
2018.01.28 19:14:52 5 : RXArduino Dispatch: U79#A5E, test gleich
2018.01.28 19:14:52 4 : RXArduino Dispatch: U79#A5E, Dropped due to short time or equal msg`

Ralf9, 2018-01-28:
Sieht so recht gut aus.
Auf das event "DMSG U79#A5E" kannst Du jetzt z.B. mit einem notify reagieren.

Du kannst mal schauen ob die Wiederholungen ausreichen. Du kannst z.B. die
"RXArduino: decoded matched MU Protocol id 79 dmsg U79#A5E length 12"
log Einträge pro Tastendruck zählen. Es sollte nie weniger als 2-3 sein.

Du kannst auch mal schauen ob das notify jedesmal reagiert, wenn Du ca 20 mal oder öfters die Klingeltaste drückst.

Falls die Wiederholungen zu wenig sind, müssen wir auch noch die Wiederholungen in den einzelnen MU-Nachrichten auswerten.

maik-dmxc, 2018-01-28:
habs grade mit nem doif getestet -> iss bloöd weil RXArduino:&DMSG:U79#A5E ständig da steht.
dadurch bekomme ich jetze ständig Telegram Nachrichten :)

werde mal ein notify probieren!

ps. danke fürs reopen

sidey79, 2018-01-28:
@maik-dmxc 
Wenn Du auf das Event und nicht aus Reading reagieren willst, dann musst Du das eher so angeben;
[RXArduino:DMSG:U79#A5E]

Grüße Sidey

maik-dmxc, 2018-01-28:
mit dem notify

> RXArduino:.* {if ($EVTPART1 eq "U79#A5E") {fhem ("set teleBot send Klingel!")};}
gehts perfekt!!!

es kommt immer nur eine Telegram Nachricht pro klingeln

:)

sidey79, 2018-01-28:
Mag funktionieren, aber im Notify reagierst Du auf alle Events von RXArduino um dann mit einer IF Abfrage den Inhalt zu prüfen.

Du kannst sowohl im Notifiy als auch DOIF den exakten Wert abfragen ohne das Zusätzliche IF.
Das Event ist eine Zeichenkette in dem alles steht und im Notify /DOIF kannst Du eine Regex eingeben, auf was Du reagieren möchtest.

maik-dmxc, 2018-01-29:
@sidey79 

hab jetze das als notify
`RXArduino:DMSG:U79#A5E {fhem ("set teleBot send Klingel!")};`

besser?

[3 more replies]