RFD-FHEM/SIGNALDuino #86: Initialisierung von MsMoveCount

sidey79, 2018-02-13:
@Ralf9 

Habe ich einen Fehler gefunden oder ist das so gewollt?

https://github.com/RFD-FHEM/SIGNALDuino/blob/8f7418eb1012dc75ee458b5d5c1a8be25e52548b/src/_micro-api/libraries/signalDecoder/src/signalDecoder.h#L90

Sollten wir msMoveCount nicht mit per default auf 3 initialisieren? Denn der einzige mir bekannte Weg ist, dass wir z.B. vollen Puffer oder etwas anderes haben (valid==false  muss erst zutreffen) .

https://github.com/RFD-FHEM/SIGNALDuino/blob/8f7418eb1012dc75ee458b5d5c1a8be25e52548b/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L157-L159

Wir  machen wir aber vorher noch zwei mal reset, weil in processMessage auf MsMoveCount >0 geprüft 
wird, gibt es kein buffermove wenn eine MS Nachricht gefunden wurde und das m_truncated flag wird dadurch auch nicht gesetzt:

https://github.com/RFD-FHEM/SIGNALDuino/blob/8f7418eb1012dc75ee458b5d5c1a8be25e52548b/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L494-L499

Am ende von ProcessMessage hauen wir den kompletten Puffer weg:
https://github.com/RFD-FHEM/SIGNALDuino/blob/8f7418eb1012dc75ee458b5d5c1a8be25e52548b/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L794-L797

Und in doDetect() passiert es zwangsweise halt noch mal, was sich nicht ohne weiteres vermeiden lässt.

https://github.com/RFD-FHEM/SIGNALDuino/blob/8f7418eb1012dc75ee458b5d5c1a8be25e52548b/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L147-L155

Es wurde also eine Nachricht ausgegeben, dann der Puffer mit allen Wiederholungen gelöscht, damit wir nun msMoveCount auf 3 setzen.
https://github.com/RFD-FHEM/SIGNALDuino/blob/8f7418eb1012dc75ee458b5d5c1a8be25e52548b/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L157-L160

Ist das ein Bug oder steckt etwas hinter dieser Methode?

Ralf9, 2018-02-14:
> MsMoveCount = 0; 

Ja, da ist `MsMoveCount = 3; ` besser. Dieser Bug hat nur evtl eine Auswirkung, wenn die erste Nachricht eine MS ist.

Ja, das mit den MS-Wiederholungen ist nicht sauber gelöst. Um dies sauber zu machen, muss in der doDetect und processMessage einiges geändert werden. U.a. muß in der processMessage vor der Ausgabe der MS-Nachricht der MsMoveCount geprüft werden.

Ich habe das mit den MS-Wiederholungen in meiner 3.3.2 besser gelöst. Dazu waren größere Änderungen in der doDetect und processMessage nötig.

sidey79, 2018-02-14:
@Ralf9 
Bin nicht so sicher, ob der Bug nicht öfters auftreten kann, da msMoveCount nicht in reset() behandelt wird.

Was deine 3.3.2 Anpassungen angehen, da komme ich leider nicht mehr mit. Das sind so viele Änderungen, die ich leider nicht nachvollziehen kann.

Mein Ziel ist es immer noch die 3.3.1 mal endlich fertig zu bekommen (MC Decoder halt noch immer) und dann mal grundlegend über ein paar Änderungen nachdenke.
Beim unittest/Integrationstest habe ich leider auch festgestellt, dass es nicht so einfach ist, die beiden Decoder Klassen ordentlich zu testen.

Der Einsatz von flags (bool Variablen) sollte auch wohl durchdacht sein. Da ergeben sich leider doch ruck zuck abhängigkeiten und der Durchblick geht verloren. Vermutlich ist es auch besser in Zukunft flags so zu benahmen, was sie auslösen sollen oder eben nicht, anstelle den Zustand zu beschreiben.

commit e00dac8f (2018-02-14): signalDecoder.cpp

msMoveCount bei jedem reset() initalisieren  fixes #86
closed 2019-02-25 by sidey79