RFD-FHEM/SIGNALDuino #90: S 300 TH - 868Mhz

HomeAutoUser, 2018-04-04:
Hallo, wie auch schon bei dem [THR128  ](https://github.com/RFD-FHEM/SIGNALDuino/issues/75#issuecomment-378677478) wird auch dieser Sensor mit dem aktuellen [DEV-R33 Commit](https://github.com/RFD-FHEM/SIGNALDuino/tree/dev-r33_cc1101) empfangen. Sobald ich einen downgradde durchführe, werden die beiden Sensoren wieder empfangen.

HomeAutoUser, 2018-04-04:
WS 7000-20 sehr auffällige Fehlermeldungen ständig
>2018.04.04 19:34:37 4: nano_433Mhz/msg READredu: MU;P0=-6328;P1=752;P2=-469;P3=271;P4=-951;D=01212121212121212121234121234123434343412343412123434123412123412341212341234121234121234123434341212343434121234121212343434343412343412341234123434123434341212;CP=3;R=0;
2018.04.04 19:34:37 4: nano_433Mhz: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2018.04.04 19:34:37 4: nano_433Mhz: WS2000 Sensortyp 4 Adr 7 - ERROR check XOR

>2018.04.04 19:37:18 4: nano_433Mhz/msg READredu: MU;P0=-10916;P1=180;P2=777;P3=-444;P4=282;P5=-933;D=0102323232323232323232345232345234545454523454545452345234523234523452323454545232345232345234545452323454545232345232323454545454523452323452345232323454545452323;CP=4;R=0;
2018.04.04 19:37:18 4: nano_433Mhz: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2018.04.04 19:37:18 4: nano_433Mhz: WS2000 Sensortyp 4 Adr 7 - ERROR check XOR

Scheinbar erkennt die FW etwas anders, das die Prüfsumme nicht mehr stimmt.

sidey79, 2018-04-04:
Hast Du zum vergleich vielleicht ein paar Meldungen von einer funktionierenden FW Version?

HomeAutoUser, 2018-04-04:
Bin auf der Suche aber bisher negativ. Der Cul empfängt diesen aber kein SIGNALduino. *forste weiter*

HomeAutoUser, 2018-04-04:
von allen 2 Typen , welche bei mir auf 3 Sensoren verteilt sind.

>2018.04.04 23:05:29 4: nano_868Mhz/msg READredu: MU;P1=846;P2=-375;P3=358;P4=-858;D=0121212121212121212123434121212343412121234123434123412341212341234121234341212123412123412341212341234343434123434121212;CP=3;R=32;
2018.04.04 23:05:29 4: nano_868Mhz: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2018.04.04 23:05:29 4: nano_868Mhz: WS2000 Sensortyp 1 Adr 1 - Thermo/Hygro
2018.04.04 23:05:29 4: nano_868Mhz: decoded matched MU Protocol id 60 dmsg K11261244 length 32 RSSI = -58

>2018.04.04 23:07:01 4: nano_868Mhz/msg READredu: MU;P1=858;P2=-362;P3=371;P4=-843;D=0121212121212121212123434121212341212121234121212343412121212341234121234341234123412121212343412341234343412343434123412;CP=3;R=246;
2018.04.04 23:07:01 4: nano_868Mhz: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2018.04.04 23:07:01 4: nano_868Mhz: WS2000 Sensortyp 1 Adr 0 - Thermo/Hygro
2018.04.04 23:07:01 4: nano_868Mhz: decoded matched MU Protocol id 60 dmsg K01085250 length 32 RSSI = -79

>2018.04.04 23:08:26 4: nano_868Mhz/msg READredu: MU;P0=-849;P1=855;P2=-370;P3=369;D=0121212121212121212123030121212303012121230123030123012301212301230121230303030123012123012301212301230301212123030121212;CP=3;R=32;
2018.04.04 23:08:26 4: nano_868Mhz: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2018.04.04 23:08:26 4: nano_868Mhz: WS2000 Sensortyp 1 Adr 1 - Thermo/Hygro
2018.04.04 23:08:26 4: nano_868Mhz: decoded matched MU Protocol id 60 dmsg K11267244 length 32 RSSI = -58

>2018.04.04 23:09:57 4: nano_433Mhz/msg READredu: MU;P0=136;P1=-947;P2=424;P3=-2168;P4=1044;P5=-467;P6=760;P7=269;D=0123456565656565656565657165657165717171716571656571657171716565716571656571716571657171657165717171656571717171657165656571717171716571657165717171717165717165717170;CP=7;R=63;
2018.04.04 23:09:57 4: nano_433Mhz: Fingerprint for MU Protocol id 60 -> WS2000 matches, trying to demodulate
2018.04.04 23:09:57 4: nano_433Mhz: WS2000 Sensortyp 4 Adr 7 - Thermo/Hygro/Baro
2018.04.04 23:09:57 4: nano_433Mhz: decoded matched MU Protocol id 60 dmsg K7434523587A7 length 48 RSSI = -42.5

HomeAutoUser, 2018-04-05:
@sidey79 ACHTUNG,
[es liegt mit dieser Änderung zusammen](https://github.com/RFD-FHEM/RFFHEM/issues/156#issuecomment-377814633). Wenn jemand in der stable sich beschwert, dann deswegen.
Änderung Rückgänging gemacht um die Warnung zu erhalten und sofort gehts!

Das muss anders gelöst werden und auf die Spur gegangen werden.

sidey79, 2018-04-05:
Danke für den Hinweis. Ich schau mir das an, an welcher Stelle es nicht passt

elektron-bbs, 2018-04-05:
@HomeAutoUser

HomeAutoUser, 2018-04-05:
@elektron-bbs 

>} until ($index >= $datalength - 1);

sieht gut aus. Sie laufen wieder.

elektron-bbs, 2018-04-05:
OK, @sidey79

elektron-bbs, 2018-04-06:
Nach einem Update, Neustart und 20 Minuten Probelauf sind erst mal keine Auffälligkeiten ersichtlich.

sidey79, 2018-04-08:
https://github.com/RFD-FHEM/RFFHEM/commit/28ff334e834bb7c5eb6d1a04905d62d39b58214b fixes #90

closed 2018-04-08 by sidey79