RFD-FHEM/SIGNALDuino #91: KEINE Eindeutige MU - MS Erkennung bei Wiederholungen eines Signales

HomeAutoUser, 2018-04-04:
Daten von dem intertechno - PIR-1000 wo zwischendrin die Erkennung wechselt. :-( MS. [Protokoll IT-v3](https://www.sweetpi.de/blog/329/ein-ueberblick-ueber-433mhz-funksteckdosen-und-deren-protokolle) mit sync

>2018.04.04 20:24:07 4: nano_433Mhz/msg READredu: MS;P0=-287;P1=256;P2=-1080;P5=-2586;D=4151012101210121210121010121210121010121210101212101012101212101012101210121210101210121012101212101210101210121210121010121012121014;CP=1;SP=5;R=17;
2018.04.04 20:24:07 4: nano_433Mhz/msg READredu: MU;P0=-1108;P1=236;P2=-270;P4=-9524;P5=-2584;P6=-356;D=0101210121210101212101012121012101012121012101210101212101210121012101012101212101210101210121210121010121415121012101210101210121210101210121210101212101012121012101012121012101210101212101210121012101016101212101210101210121210121010121415121012101210;CP=1;R=16;
2018.04.04 20:24:07 4: nano_433Mhz/msg READredu: MU;P0=-1083;P1=257;P2=-284;P4=-9522;P5=-2588;D=0121012121010121012121010121210101212101210101212101210121010121210121012101210101210121210121010121012121012101012141512101210121010121012121010121012121010121210101212101210101212101210121010121210121012101210101210121210121010121012121012101012141512;CP=1;R=15;
2018.04.04 20:24:07 3: radino_433Mhz IT: IT_00000000000000000000000000000000000000000000000000000 dimup->dimup
2018.04.04 20:24:07 4: nano_433Mhz/msg READredu: MU;P0=-1114;P1=237;P2=-289;P4=-9528;P5=-2576;D=0101012121012101212101012101212101012121010121210121010121210121012101012121012101210121010121012121012101012101212101210101214151210121012101012101212101012101212101012121010121210121010121210121012101012121012101210121010121012121012101012101212101210;CP=1;R=15;
2018.04.04 20:24:07 3: radino_433Mhz IT: IT_V3_da910c9 an->on
2018.04.04 20:24:08 4: nano_433Mhz/msg READredu: MS;P0=-1078;P1=238;P2=-296;P5=-2574;D=3151210121012101012101212101012101212101012121010121210121010121210121012101012121012101210121010121012121012101012101212101210101213;CP=1;SP=5;R=16;m=2;
2018.04.04 20:24:08 4: nano_433Mhz/msg READredu: MS;P0=-1092;P1=242;P2=-291;P5=-2573;

HomeAutoUser, 2018-05-04:
Hallo @sidey79,
passend zu dem MU - MS Problem...

Ich versuche mich gerade an der Einbindungen von diversen Signalen.
Dabei ist mir aufgefallen, das die MS Ausgabe nicht mit P0 beginnt. Die Angabe des P0 oder teilweise P1 fehlt bei der Ausgabe. Kannst du dir erklären wieso der Decoder dies verschluckt?

>MS;P2=806;P3=-1719;P4=-917;P5=1576;P6=-6670; D=26232324532323232323232324532324545323232323232453;CP=2;SP=6;R=96;m=2;
MS;P2=806;P3=-1719;P4=-917;P5=1576;P6=-6670;P7=-32001;D=2623232453232323232323232453232454532323232323245327772;CP=2;SP=6;R=96;
MS;P3=793;P4=-1723;P5=-916;P6=1576;P7=-6650;			D=37343435643434343434343435643435656434343434343564;CP=3;SP=7;R=95;m=0;
MS;P1=-6609;P2=805;P3=-1721;P4=-917;P5=1578;D=21232324532323232323232324532324545323232323232453;CP=2;SP=1;R=94;m=2;
MS;P1=-6609;P2=805;P3=-1721;P4=-917;P5=1578;D=21232324532323232323232324532324545323232323232453;CP=2;SP=1;R=94;m=1;
MS;P1=-6609;P2=805;P3=-1721;P4=-917;P5=1578;D=21232324532323232323232324532324545323232323232453;CP=2;SP=1;R=94;m=0;

HomeAutoUser, 2018-05-04:
> Wenn P0 ein Wert ist, der nicht in D enthalten ist, dann wird P0 auch nicht ausgegeben.
D wird bei MS Nachrichten ja auf den Bereich reduziert, der mit einer Sync Sequenz beginnt und endet.

Ist das Ganze auch darauf zu beziehen wenn P1 oder P2 fehlt?

sidey79, 2018-09-12:
@homeautouser

Ist das Problem noch vorhanden?

HomeAutoUser, 2018-09-13:
@sidey79
Schaue ich Anfang nächster Woche, bin mal im verlängerten Wochenende ;)