RFD-FHEM/SIGNALDuino #92: IT Senden - aktuelle DEV

HomeAutoUser, 2018-04-04:
nun 2 mal probiert .... aktuelle DEV: V 3.3.1-RC4_dev-r33_cc1101_180312 SIGNALduino - compiled at Apr 4 2018 19:06:20 auf dem Empfänger und senden zum IT geht nicht mehr.

DEV SIGNALDuino-dev-r33_cc1101_180304 auf den Empfänger und sofort gehts wieder.
Es wurden nur Änderungen der FW vorgenommen. 2x reproduziert :-( *wegduck*

HomeAutoUser, 2018-04-04:
Wenn ich eine andere FW auf den 2. Empfänger mache, so kann ich via IODEV umstellen das ganze Verhalten auch nachstellen. Einmal sehen mit der SIGNALDuino-dev-r33_cc1101_180304 geht und auf dem anderen Empfänger wo 3.3.1-RC4_dev-r33_cc1101_180312 SIGNALduino drauf ist, geht die Birne nicht mehr aus.

sidey79, 2018-04-04:
Hast Du das fertige binary genutzt oder selbst compiliert und eventuell irgendwas an dem code verändert?

HomeAutoUser, 2018-04-04:
Ich habe es selbst kompiliert ohne was am Code zu ändern.

sidey79, 2018-04-05:
Der f3b281e6473f85f6346d2cd45b385e318ddf6394 commit ist doch der aktuellste. Wenn es damit geht, dann sind das die aktuellen Quellen.

HomeAutoUser, 2018-04-05:
>Der f3b281e commit ist doch der aktuellste. Wenn es damit geht, dann sind das die aktuellen Quellen.

Was definierst du als aktuellste? Wenn ich in die Liste sehe, dann sehe ich noch den 3dd6473 | e70cfb6 | 4819ec4 | 47a3c40 | 080321c | b9e4abc | .... bis hin zu dem e791538 welcher für mich der aktuellste wäre. ;-)

Zwischen dem f3b281e 4.März und dem letzte e791538 12. März liegen 12 Änderungen. Wieso geht es bei dem letzten e791538 , also aktuellen dann nicht?

Grund des Tests, war, dir zu zeigen, das sein 12 Änderungen das IT nicht mehr geht zu senden.
Ich sehe dies so, das bei jeder Änderung welche du commitest, der Code ja funktionieren muss, ODER sehe ich das VERKEHRT?

** oder schreiben wir beide aneinader vorbei **

sidey79, 2018-04-05:
Sorry, hatte mich da blenden lassen... Wenn man den Commit anklickt, dann gibt es natürlich in der Historie nichts neueres....

Also mit Commit 3dd6473a8cb498b1330e950b618282ec6e1123e6 geht es nicht mehr. Das können wir wohl festhalten.

Mit cc1101 oder ohne? Welche Variante ist betroffen?

HomeAutoUser, 2018-04-05:
>Also mit Commit 3dd6473 geht es nicht mehr. Das können wir wohl festhalten.

Richtig!

>Mit cc1101 oder ohne? Welche Variante ist betroffen?

mit einem CC1101

sidey79, 2018-04-05:
https://github.com/RFD-FHEM/SIGNALDuino/blob/e791538a4d26048a1f06bf6d35f4bfba8b9c7f77/RF_Receiver/RF_Receiver.ino#L37

war aktiv oder?

sidey79, 2018-04-05:
Habe jetzt auch noch mal die SIGNALDuino_nanocc1101.hex (
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.3.1-RC4) auf einen Nano mit cc1101 geflasht.

Senden klappt. Eine ITv3 Funksteckdose lies sich problemlos schalten.
Ich habe beim Testen aber festgestellt, dass meine ITv1 clon Dosen nicht mehr empfangen werden.

HomeAutoUser, 2018-04-13:
>Ich habe beim Testen aber festgestellt, dass meine ITv1 clon Dosen nicht mehr empfangen werden.

gibt es hier neue Erkenntnisse?

closed 2018-04-16 by HomeAutoUser