RFD-FHEM/SIGNALDuino #94: 3.3.1 - RC 4 für Arduino nano (ATMEGA328P) mit CC1101 kompilieren

plin2, 2018-04-14:
Hi, mit welchen Options muss ich den RF_Receiver in der Arduino IDE kompilieren um ihn auf einem nano mit cc1011 einsetzen zu können?

Hintergrund der Frage: Wenn ich die SIGNALDuino_nanocc1101.hex flashe kann ich mit dem cc1101 kommunizieren und Datenpakete empfangen. Die selbstkompilierte Version mit 
#define ARDUINO_ATMEGA328P_MINICUL 1
redet zwar mit dem cc1101, ich empfange aber nichts.

elektron-bbs, 2018-04-14:
Beim MiniCUL sind Send- und Receive-Pins vertauscht. Du kannst also nichts empfangen.
So sollte es funktionieren:

// Config flags for compiling correct options / boards Define only one
//#define ARDUINO_ATMEGA328P_MINICUL 1
//#define ARDUINO_AVR_ICT_BOARDS_ICT_BOARDS_AVR_RADINOCC1101 1;
#define OTHER_BOARD_WITH_CC1101  1

sidey79, 2018-04-14:
Welches Board hast Du denn?
Einen minicul oder einen nano.

Vorcompilierte Versionen gibt es für beide.
Wenn was vertauscht ist, passe ich es natürlich an.

elektron-bbs, 2018-04-14:
Die Zuordnung der Pins in der Firmware passt schon. Die Send- und Receive-Pins sind hardwaremäßig beim MiniCUL vertauscht.

sidey79, 2018-04-14:
Verstehe ich jetzt nicht ganz.

Die pins sollten in der Firmware eigentlich passend zugeordnet werden.

Mich irritiert halt, dass Du minicul schreibst, aber eine Firmware für einen nano verwendet hast.

plin2, 2018-04-14:
@Sidey79: Ich habe einen Arduino nano atmega328p und einem cc1101 (SIGNALduino wie im FHEM Wiki beschrieben). Die Optionen im RF_Receiver.ino haben keine Kommentare was wofür zu verwenden ist. Eine Doku der Optionen habe ich unter "SIGNALduino selbst compilieren" auch nicht gefunden, also habe ich dem Code entnommen:
ARDUINO_ATMEGA328P_MINICUL führt zu #define CMP_CC1101
Das klang für mich passender als OTHER_BOARD_WITH_CC1101. Dass die send/receive-PINs getauscht sind ist mir sehr wohl aufgefallen, aber ich dachte mit dem ARDUINO_ATMEGA328P_MINICUL würde ich richtig liegen.

sidey79, 2018-04-14:
Das stimmt. Die Compiler Optionen sind nicht dokumentiert.

OTHER_BOARD_WITH_CC1101
Passt für einen Nano.

Was hat mit der vorcompilierte Version nun nicht funktioniert? Das irritiert mich ja doch ein wenig.

plin2, 2018-04-14:
>Was hat mit der vorcompilierte Version nun nicht funktioniert? Das irritiert mich ja doch ein wenig.
mit der ist alles in Ordnung. Ich nähere ich langsam dem 2-FSK-Protokoll (Thema "unbekannte Funkrptokolle" im FHEM Forum). Der für mich einfachste Ansatz war zunächst mal einen funktionsfähigen SIGNALduino zu haben, dann sukzessive die cc1101-Register anzupassen bis ich bei 868.300 MHz, Bandbreite 58 kHz, Deviation 23 kHz und 2-FSK etwas empfange und das verstehe bzw. wiedererkenne. Senden wäre dann der nächste Schritt.

sidey79, 2018-09-12:
@plin2

Ist das Problem gelöst? Wenn ja dann schließe doch bitte diesen Fall.

closed 2018-09-17 by plin2