RFD-FHEM/SIGNALDuino #95: getUptime gibt keine Werte größer 49 Tage und 17 Stunden zurück

Ralf9, 2018-04-30:
Mit der getUptime() in der RF_Receiver.ino scheint irgendwas nicht zu passen.

Ich habe gestern Abend bei einem get uptime folgendes erhalten:
`49 10:19:33 |2018-04-29 22:33:00`
und heute folgendes:
`0     15:53:37  |  2018-04-30 21:10:00`

Dies müsste doch ein "unsigned long" sein?
`seconds += (( long(4294967295) / 1000 )*times_rolled);`

```
inline unsigned long getUptime()
{
	unsigned long now = millis();
	static uint16_t times_rolled = 0;
	static unsigned long last = 0;
	// If this run is less than the last the counter rolled
	unsigned long seconds = now / 1000;
	if (now < last) {
		times_rolled++;
		seconds += (( long(4294967295) / 1000 )*times_rolled);
	}
	last = now;
	return seconds;
}
```

sidey79, 2018-05-07:
fixed with cf4f54c97e11e0f032b359030fad9908766b2593

elektron-bbs, 2019-06-20:
Wenn das Problem erledigt ist, könnte das Issue geschlossen werden.