RFD-FHEM/SIGNALDuino #98: Arduino hang up - no receive - must restart

HomeAutoUser, 2018-09-23:
Erscheinung:
Bei diversen SIGNAL Kombinationen bzw. bestimmten Empfängen der Signale nacheinader oder bei 
bestimmten Konstellationen der Empfänge hängt sich der SIGNALduino auf und die L-Led blink. Kein Empfang ist merh möglich.

Ausgeschlossen:
Verschiedene Hardware Arduino´s

Rekonstruktion:
- "vermutlich" bestimme Aufeinadertreffen vieler Signale
-  an 2 verschiedenen Orten bisher mit unterschiedlicher Hardware aufgetreten

Hardware:
- aktuelle DEV-r33 cc1101 Version

sidey79, 2018-09-23:
Hatte ich eine Zeit lang auch Mal. Jetzt irgendwie nicht mehr.
Kann es sein, dass es beim Senden passiert?

elektron-bbs, 2018-09-23:
Ich hatte den Effekt letztes Wochenende auch erstmalig.
Von der Firmware wird doch eigentlich die interne LED nicht angesteuert? Diese ist am PB5 / SCK angeschlossen.
Wenn die Firmware versagen würde, müsste ja eigentlich der Watchdog greifen.

HomeAutoUser, 2018-09-30:
Am gestrigen Tage musste ich feststellen, das beim Senior ebenso der Fehler auftrat.
Nach einem Restart lief das Device dann 10min bis zum erneuten Fehler. Ein Restart und nun läuft er erstmal wieder.

Einzige Parallele seine Umgebung zu meiner Umgebung = sehr viele Sender senden und somit viele Signale auf einmal. Die Vermutung könnte nahe liegen, das bei einem bestimmten Aufeinandertreffen etwas zum ERROR führt.

Wenn es nicht die FW ist, kann es sein, das wir den Arduino zu sehr "stressen"? Bei  meinem radino war das Fehlerbild bisher nicht aufgetreten.

sidey79, 2018-09-30:
Klingt so wie diese Beschreibung oder?

https://arduino.stackexchange.com/questions/18369/reset-arduino-uno-r3-from-serial-when-it-freezes-stops-looping

HomeAutoUser, 2018-09-30:
>Klingt so wie diese Beschreibung oder?

genau so in etwa. Leider habe ich bisher nichts finden können geschrieben, das es beim radino funktiniert. Mit den Tests wie ich bisher gefahren ware, klappte es nicht.

sidey79, 2018-10-01:
Als Workaround könnte wir in das Perl Modul ein toggle der dtr Leitung einbauen.

Müsste aber verifiziert werden, ob das hilft.
$PortObj->pulse_dtr_on($milliseconds);
$PortObj->pulse_dtr_off($milliseconds);

sidey79, 2018-10-01:
Mal schauen ob es so funktioniert:
https://github.com/RFD-FHEM/RFFHEM/commit/6fcbde9e09010b494d396fa2afe1cd1312ec9cb1

sidey79, 2018-10-03:
Ich habe mich auch dran gesetzt die String Klasse durch C Strings zu ersetzen.
Dabei habe ich viele Funktionen in eigene Header verschoben, damit ich das ganze auch Mal mit der ESP Version verheiraten kann.

https://github.com/RFD-FHEM/SIGNALDuino/commits/dev-r33_cc1101-serial-parse

sidey79, 2018-10-03:
Ich habe auch überlegt, ob es vielleicht so ist, dass der Watchdog nicht zurück gesetzt wird, weil die Serielle Verbindung deaktiviert wurde.

Hier eine Variante, bei der das nicht passiert.

d475554

Alternativ könnte man den Watchdog in der Setup Routine auch deaktivieren.

sidey79, 2018-10-04:
@HomeAutoUser 

Ich habe so einen Fall reproduzieren können.

Wenn der Watchdog den Arduino resettet, dann reagiert er anschließend nicht mehr auf serielle verbindungsversuche.
Ob DTR Toggeln einen sauberen reset macht bin ich nicht sicher. Habe es manuell probiert aber nicht geschafft.

So ähnliches Verhalten habe ich schon bei anderen Projekten gehabt aber immer auf andere Umstände geschoben.

Eigentlich habe ich den Watchdog eingebaut, damit der Microcontroller wieder in einem definierten Zustand kommt. Workaround wäre ihn zu deaktivieren.

sidey79, 2018-10-04:
https://github.com/arduino/Arduino/issues/4492

sieht für mich so aus, als ob wir da wenig machen können

HomeAutoUser, 2018-10-04:
Wenn wir dem hilflos ausgesetzt sind, bekommen wir den Arduino dazu, mit einem FHEM reset beim Device, die Spannung kurz zu unterbrechen, das man nicht immer gezwungen ist, manuell USB an/ab zu machen?

elektron-bbs, 2018-12-18:
Diese Nacht hatte sich mein SIGNALduino wieder einmal "aufgehangen". Der Reset aus FHEM scheint nichts zu bringen. Ich musste ihn heute morgen kurz von der Betriebsspannung trennen, dann lief er wieder. Der Reset-Taster ist leider nicht von außen zugänglich. Das Gehäuse aufzuschrauben, um diesen zu betätigen, war mir heute morgen zu umständlich. Hier das Log dazu:
[fhem_2018-12-18.log](https://github.com/RFD-FHEM/SIGNALDuino/files/2691410/fhem_2018-12-18.log)

elektron-bbs, 2018-12-23:
... und schon wieder. Ich werde dem Teil wohl einen externen Reset-Taster spendieren müssen...
[fhem_2018-12-23.log](https://github.com/RFD-FHEM/SIGNALDuino/files/2706297/fhem_2018-12-23.log)

sidey79, 2018-12-23:
Finde ich echt spannend, hast Du eine Idee wie wir herausfinden könnten, an welcher Stelle sich der Arduino aufhängt?

elektron-bbs, 2018-12-23:
Nein, absolut nicht. Ich versuche jetzt ungenutzte Ports auf einen definierten Pegel zu legen (Pullup ein).

elektron-bbs, 2019-01-10:
Die Pullups einzuschalten hat auch nichts gebracht :-(
Ich werde dann mal deine neue Version ausprobieren.
[fhem_2019-01-09.log](https://github.com/RFD-FHEM/SIGNALDuino/files/2746771/fhem_2019-01-09.log)

HomeAutoUser, 2020-01-13:
@elektron-bbs 
ich denke der Fall kann geschlossen werden oder wie ist das aktuelle Verhalten bei dir?

elektron-bbs, 2020-01-13:
Das er sich ab und an "aufhängt" ist zwar nicht erledigt, aber zumindest funktioniert der Reset jetzt von FHEM aus.

hackfred64, 2020-12-18:
Ich denke ich habe das gleiche Problem, habe nach dem "abarbeiten" dieses Threads aber noch nicht verstanden wie das mit dem Reset von FHEM aus funktionieren soll. Könnt ihr mich da bitte noch einmal "aufgleisen"?

Danke!

HomeAutoUser, 2020-12-20:
Hallo @hackfred64,
gern helfen wir dir. Kannst du bitte mehr Input liefern.
- welche genaue Hardwarekonstelation
- welche Firmware
- von welchen Zeitraum reden wir bei den Aufhängern

hackfred64, 2020-12-21:
Nun, ich habe einen zugegebenermaßen etwas nerdigen Aufbau. Vor etwas mehr als einem halben Jahr hab ich mir Rolladen mit Somfy RTS Fernbedienung installiert. Da ich ohnehin FHEM für andere Aufgaben/Protokolle schon in Benutzung hatte war ich neugierig geworden, da ich mit dem Signalduino offenbar auch Somfy RTS ansprechen kann.
Wie das bei mir so ist, ich gehe dann in meinen Keller und schau mal was sich so findet um probieren zu können ;-)
Es fand sich noch ein Raspi Zero, ein CC1101 und ein altes Arduino Serial Board sowie einige Atmega328! Nicht schlecht für einen Samstag Nachmittag an einem verregneten Wochenende.
Kurz zusammengefasst: damit konnte ich starten! Mit dem Serial Board konnte ich den Arduino direkt mit der seriellen Schnittstelle des Raspi verbinden, mir also ein USB-Serial Interface sparen. Nachdem die Rolladen aus FHEM steuerbar waren wurde ich mutiger: da sollten sich doch auch Obi Steckdosen (GT-9000) ansteuern lassen. Prima! Auch das funktioniert.
Nur mit der Einschränkung der Abstürze des Signalduino der sich nach einigen wenigen Tagen, bis zu mehreren Wochen, aufhängt. Heilen kann ich den Zustand durch Neustart des Signalduino (Power Cycle) sowie "reopen" von FHEM aus.
Meine installierte Firmware: nanocc1101R3.3.1-RC10
Durch meine direkte serielle Anbindung kann ich natürlich z.B. keinen DTR Pin vom USB Interface für einen Reset missbrauchen. Da müsste ich mir einen Patch bauen mit dem ich einen der GPIOs dafür nutzen könnte und auf den Reset des Arduino legen.

Schonmal Danke dafür wenn ihr mir hier ein wenig mehr Insights geben könntet. Dann kann ich über die Feiertage weiter experimentieren :-).

/ Stefan

elektron-bbs, 2020-12-21:
> Meine installierte Firmware: nanocc1101R3.3.1-RC10

Zuerst würde ich es mal mit einer aktuellen Firmware-Version versuchen. Vielleicht erledigt sich das Problem dadurch ja schon selbst.

Ich habe hier auch den Effekt, das sich die Arduino Nano unterschiedlich verhalten, je nach Typ des Serial-USB-Wandlers.
Arduinos mit FTDI werden mit einem Reset aus FHEM zuverlässig neu gestartet. Bei einem Arduino mit CH340 funktioniert das nicht.

hackfred64, 2020-12-21:
> Wahrscheinlich funktioniert das mit direkter serieller Anbindung dann auch nicht. Lässt sich denn der Arduino direkt aus FHEM flashen?

Wahrscheinlich (ohne Bootloader?!) nicht. Ich meine ich hätte damals direkt das Hexfile geflasht. Da könnte natürlich der Bootloader dabei gewesen sein. Ich hatte aber zumindest den Versuch gemacht die Sourcen zu compilieren, bin mir aber nicht mehr sicher ob ich mit denen dann ohne Bootloader geflasht habe. Ich besorge mir jetzt erst einmal das Hex File der aktuellen Firmware und werde berichten!

/ Stefan

hackfred64, 2020-12-21:
Übrigens, vorab noch eine Beobachtung:
Das Ganze tritt bei mir im Sommer deutlich seltener auf. Hört sich absurd an, aber: anders als im Sommer schalte ich im Winter regelmäßig die Steckdosen (Licht). Das könnte die Vermutungen am Anfang des Threads unterstützen, dass etwas beim Senden passiert...

hackfred64, 2020-12-21:
So, neue Firmware ist drauf (über ISP, aus FHEM heraus klappt wie erwartet nicht): SIGNALDuino_nanocc11013.4.0.hex
Ich werde berichten ob sich das Verhalten dadurch verändert.
Bis dahin, Danke!

hackfred64, 2021-01-11:
So, finales Feedback: seit 21 Tagen reibungsloser Betrieb, keine Ausfälle mehr nach dem Firmware Update. Danke an das Team! Ich denke der Thread kann somit wieder geschlossen werden.

closed 2021-06-08 by HomeAutoUser