RFD-FHEM/SIGNALDuino #105: Message so not possible, following 2 negative pulses

HomeAutoUser, 2018-12-30:
## Expected Behavior
- message so not possible
- negative than negative does not work

## Actual Behavior
- message is issued with 2 negative in a row
`2018.12.29 00:01:45 4: sduino_433/msg READredu: MU;P0=495;P1=-961;P2=-1955;P3=-3907;P4=136;P5=-32001;P6=360;P7=-428;D=03010201010201010102010102010101010101020102010102020202020102010202010201455670;CP=0;R=14;`

## Steps to Reproduce the Problem

  1. reception with current DEV in the strong transmitter jungle
  2.
  3.

## Specifications

  - Microcontroller: Arduino Nano
  - Version (Firmware): V 3.3.1-RC10 SIGNALduino cc1101 - compiled at 09 November 2018 01:43:10

  - Versionmodul (FHEM Module): v3.3.3-dev_19.12.

sidey79, 2018-12-30:
Seltsames verhalten. Es ist aber nicht das einzigste wo ich zwei gleiche Pulse in den Nachrichten gesehen habe.
Sehr seltsam.

HomeAutoUser, 2019-01-01:
Der Fall ist auf jedenfall wiederkehrend.

`2019.01.01 00:00:13 4: sduino_433/msg READredu: MU;P0=444;P1=-1055;P2=-2063;P3=-4099;P4=-32001;D=0302010201020202010201010101010101010201020101020202020202010202010101010440;CP=0;R=211;`

HomeAutoUser, 2019-01-01:
Meine Vermutung ist, es hängt mit einer MS Verarbeitung zusammen das solcher "Müll" herauskommt.
Das ganze kann ich wiederkehrend finden.

```
2019.01.01 00:06:28 4: sduino_433/msg READredu: MS;P2=-3899;P3=485;P4=-975;P5=-1942;D=32343534343434343535343434343434343435353534353534353535353435343434353435;CP=3;SP=2;R=32;O;m2;
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 1 -> Conrad RSL v1
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:28 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#418076F45 length 36 RSSI = -58
2019.01.01 00:06:28 4: sduino_433: SD_WS07_Parse SD_WS07 (P7#418076F45) length: 9
2019.01.01 00:06:28 4: sduino_433: SD_WS07_Parse SD_WS07 converted to bits 01000001 1000 000001110110 1111 01000101
2019.01.01 00:06:28 4: sduino_433: using longid SD_WS07_TH model SD_WS07_TH
2019.01.01 00:06:28 4: sduino_433: SD_WS07_TH_411 id=41, channel=1, temp=11.8, hum=69, bat=ok
2019.01.01 00:06:28 4: sduino_433/msg READredu: MS;P2=-3904;P3=493;P4=-979;P5=-1960;D=32343534343434343535343434343434343435353534353534353535353435343434353435;CP=3;SP=2;R=32;O;m1;
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 1 -> Conrad RSL v1
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:28 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#418076F45 length 36 RSSI = -58
2019.01.01 00:06:28 4: sduino_433 Dispatch: P7#418076F45, Dropped due to short time or equal msg
2019.01.01 00:06:28 4: sduino_433/msg READredu: MS;P2=-3902;P3=488;P4=-978;P5=-1959;D=32343534343434343535343434343434343435353534353534353535353435343434353435;CP=3;SP=2;R=32;O;m0;
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 1 -> Conrad RSL v1
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:28 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#418076F45 length 36 RSSI = -58
2019.01.01 00:06:28 4: sduino_433 Dispatch: P7#418076F45, Dropped due to short time or equal msg
2019.01.01 00:06:28 4: sduino_433/msg READredu: MS;P2=-3913;P3=497;P4=-967;P5=-1948;D=32343534343434343535343434343434343435353534353534353535353435343434353435;CP=3;SP=2;R=32;O;
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 1 -> Conrad RSL v1
2019.01.01 00:06:28 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:28 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#418076F45 length 36 RSSI = -58
2019.01.01 00:06:28 4: sduino_433 Dispatch: P7#418076F45, Dropped due to short time or equal msg
2019.01.01 00:06:29 4: sduino_433/msg READredu: MS;P0=493;P1=-979;P2=-1949;P3=-3919;D=03010201010101010202010101010101010102020201020201020202020102010101020102;CP=0;SP=3;R=0;O;m2;
2019.01.01 00:06:29 4: sduino_433: Matched MS Protocol id 1 -> Conrad RSL v1
2019.01.01 00:06:29 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:29 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#418076F45 length 36 RSSI = -74
2019.01.01 00:06:29 4: sduino_433 Dispatch: P7#418076F45, Dropped due to short time or equal msg
2019.01.01 00:06:30 4: sduino_433/msg READredu: MS;P0=500;P1=-977;P2=-1941;P3=-3917;D=03010201010101010202010101010101010102020201020201020202020102010101020102;CP=0;SP=3;R=0;m1;
2019.01.01 00:06:30 4: sduino_433: Matched MS Protocol id 1 -> Conrad RSL v1
2019.01.01 00:06:30 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:30 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#418076F45 length 36 RSSI = -74
2019.01.01 00:06:30 4: sduino_433 Dispatch: P7#418076F45, Dropped due to short time or equal msg
2019.01.01 00:06:30 4: sduino_433/msg READredu: MS;P0=500;P1=-977;P2=-1941;P3=-3917;D=03010201010101010202010101010101010102020201020201020202020102010101020102;CP=0;SP=3;R=0;m0;
2019.01.01 00:06:30 4: sduino_433: Matched MS Protocol id 1 -> Conrad RSL v1
2019.01.01 00:06:30 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:30 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#418076F45 length 36 RSSI = -74
2019.01.01 00:06:30 4: sduino_433 Dispatch: P7#418076F45, Dropped due to short time or equal msg
2019.01.01 00:06:30 4: sduino_433/msg READredu: MU;P0=500;P1=-977;P2=-1941;P3=-3917;P4=132;P5=-32001;D=030102010101010102020101010101010101020202010202010202020201020101010201024550;CP=0;R=0;
2019.01.01 00:06:30 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2019.01.01 00:06:31 4: sduino_433/msg READredu: MS;P0=-9168;P2=606;P3=-4171;P4=-2107;D=2023242423242424232423232424242424242424242423242423242424242424242324232424;CP=2;SP=0;R=216;O;m2;

2019.01.01 00:06:51 4: sduino_433/msg READredu: MS;P0=-4103;P1=460;P2=-2067;P3=-1042;D=10121312131212121312131313131313131312131213121313121212121312131212121212;CP=1;SP=0;R=212;O;m1;
2019.01.01 00:06:51 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:51 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#AE8054F5F length 36 RSSI = -96
2019.01.01 00:06:51 4: sduino_433 Dispatch: P7#AE8054F5F, Dropped due to short time or equal msg
2019.01.01 00:06:51 4: sduino_433/msg READredu: MS;P0=-4105;P1=435;P2=-2059;P3=-1062;D=10121312131212121312131313131313131312131213121313121212121312131212121212;CP=1;SP=0;R=212;O;m0;
2019.01.01 00:06:51 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:51 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#AE8054F5F length 36 RSSI = -96
2019.01.01 00:06:51 4: sduino_433 Dispatch: P7#AE8054F5F, Dropped due to short time or equal msg
[... cut ...]
2019.01.01 00:06:52 4: sduino_433/msg READredu: MS;P0=442;P1=-1051;P2=-2074;P3=-4106;D=03020102010202020102010101010101010102010201020101020202020102010202020202;CP=0;SP=3;R=209;m0;
2019.01.01 00:06:52 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.01 00:06:52 4: sduino_433: Decoded MS Protocol id 7 dmsg P7#AE8054F5F length 36 RSSI = -97.5
2019.01.01 00:06:52 4: sduino_433 Dispatch: P7#AE8054F5F, Dropped due to short time or equal msg
2019.01.01 00:06:52 4: sduino_433/msg READredu: MU;P0=442;P1=-1051;P2=-2074;P3=-4106;P4=-32001;D=0302010201020202010201010101010101010201020102010102020202010201020202020440;CP=0;R=209;
```

sidey79, 2019-01-03:
Spannend wäre zu prüfen, ob es z.B. durch die Message Reduktion falsch übergeben wird oder ob es ggf auch erst in Fhem zu einer falschen Auswertung kommt.

Das Hinzufügen der Pulse mit gleichem Vorzeichen sollte nicht funktionieren.

HomeAutoUser, 2019-01-03:
Ich werde mal die Message Reduktion ausschalten und schauen wie das Verhalten ist.

HomeAutoUser, 2019-01-03:
Ich habe mal den Versuch gestartet und kann ausschließen das es an der Message Reduktion liegt.

```
2019.01.03 17:21:53 4: sduino_433/msg READredu: MU;P0=465;P1=-1040;P2=-2042;P3=-4093;P4=-32001;D=0302010201020202010201010101010101010101010202010202020202010201010201020440;CP=0;R=213;

2019.01.03 17:22:37 4: set sduino_433 raw CDR
2019.01.03 17:22:38 4: sduino_433/HandleWriteQueue: nothing to send, stopping timer

2019.01.03 17:22:50 4: sduino_433/msg READ: MU;P0=458;P1=-1036;P2=-2052;P3=-4099;P4=-32001;D=030201020102020201020101010101010101010101020201020202020201020101020201044;CP=0;R=213;
2019.01.03 17:23:35 4: sduino_433/msg READ: MU;P0=499;P1=-1001;P2=-2008;P3=-4054;P4=-32001;D=030101010102020202020101010101010102010202010102020202020201010202010202044;CP=0;R=20;
2019.01.03 17:23:36 4: sduino_433/msg READ: MU;P0=493;P1=-982;P2=-1944;P3=-3904;P4=112;P5=-32001;D=03010201010101010202010101010101010102010101020202020202020101020202010102455;CP=0;R=34;
```

@sidey79 um auf der FHEM Seite zu schauen ggf im Modul, fehlen mir die Kenntnisse.

@Ralf9 hast du vielleicht eine Idee?

sidey79, 2019-01-03:
Schade, wenn es ohne Message Reduktion so übergeben wird, dann liegt der Fehler im Microcontroller.

Wie lange musst Du in etwa auf diesen Fehler warten?

HomeAutoUser, 2019-01-03:
Das geht fix, da er meist aller ca. 3-10minuten Auftritt wenn nicht schon eher.

sidey79, 2019-01-03:
Meine Idee wäre, ich baue eine spezielle Firmware die nur dann etwas ausgibt, wenn dieser Fehler Auftritt.
Es scheint ja fast immer die letzten beiden Stellen des Message zu treffen .

HomeAutoUser, 2019-01-03:
Das können wir so machen. Wenn du sie fertig hast, könnte ich ggf direkt aus dem Web flashen weil ich so den Sduino von der Ferne beobachten und bedienen kann.

sidey79, 2019-01-03:
Ich hab da eine Idee. Ich baue gleich eine test Firmware

sidey79, 2019-01-03:
hmm aus gleich wurden Stunden, da irgendwas mit dem automatischen deployen eines releases nicht mehr geklappt hat:

Musste da erst mal was abändern. Hier jetzt mal der Build mit einer Zusätzlichen Debug Ausgabe für MU Nachrichten (ohne MRED) 

https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.3.1-RC-nightly

HomeAutoUser, 2019-01-04:
@sidey79 vielen Dank. Ich kenne das, das aus schnell mal Stunden werden :-D

Hier ist die Erkenntnis welche nicht lange auf sich warten lies. Bei manchen "Müll" RAWNSG ist ein E & P hinzugekommen aber bei manchen auch nichts nach dem flashen.

```
2019.01.04 14:47:07 4: sduino_433/msg READredu: MU;P0=503;P1=-971;P2=-1940;P3=-3902;P4=132;P5=-32001;D=030102010102010101020101020101010101010102010201020202020201020102010201024550;CP=0;R=14;E;P;
2019.01.04 14:47:07 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate

2019.01.04 14:49:21 4: sduino_433/msg READredu: MU;P0=491;P1=-970;P2=-1953;P3=-3906;P4=132;P5=-32001;D=030102010102010101020101020101010101010102010201020202020201020102010201024550;CP=0;R=15;E;P;
2019.01.04 14:49:21 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate

2019.01.04 14:49:34 4: sduino_433/msg READredu: MU;P0=480;P1=-1019;P2=-2016;P3=-4056;P4=-32001;D=0301010101020202020201010101010101020102020101020202020202010102020201010440;CP=0;R=17;
2019.01.04 14:49:34 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate

2019.01.04 14:49:41 4: sduino_433/msg READredu: MU;P0=483;P1=-973;P2=-1965;P3=-3913;P4=136;P5=-32001;D=030102010101010102020101010101010101020101020201020202020201010202020201024550;CP=0;R=32;E;P;
2019.01.04 14:49:41 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate

2019.01.04 14:50:05 4: sduino_433/msg READredu: MU;P0=452;P1=-1053;P2=-2051;P3=-4108;P4=-32001;D=0302010201020202010201010101010101010101020202010102020202010201020201010440;CP=0;R=212;
2019.01.04 14:50:05 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
```

sidey79, 2019-01-04:
E und P sind die zwei letzen Byte aus dem Speicher:
```
E ^= 0100 0101 ^=  4  5
P ^= 0101 0000 ^=  5  0
```
Im Speicher befindet sich demnach wirklich die Sequenz 4550 die auch ausgegeben wird.

Bei den Meldungen, bei denen nichts geliefert wird, liegt noch ein anderes Problem vor. Da existiert das letzte und vorletzte Byte (zumindest da wo es sein sollte) scheinbar nicht im Puffer.

Da müssen wir noch ein wenig weiter debuggen. Schade dass ich diesen Fehler sehr selten hier habe. Da kann ich stunden warten und finde ihn nicht.

sidey79, 2019-01-04:
Ja, ich muss mir das noch mal überlegen wie ich das weiter debugge. Am besten ich prüfe nachdem ein Wert hinzugefügt wurde ob 2x das gleiche im Puffer ist. Wäre es ein Problem für dich, wenn die Auswertung in FHEM durch das debuggung "verhindert" wird?

Ralf9, 2019-01-04:
Mich würde interessieren ob der Fehler nur in der 3.3.1-RC auftritt, oder auch in meiner 3.3.2

HomeAutoUser, 2019-01-04:
@Ralf9 gern teste ich auch für dich deinige FW.
Was ist bei dir die aktuelle für den nano cc1101? Da sind so viele und auch noch Monate zurück...
Gibt es die Möglichkeit diese herunterzuladen ohne das ich sie selbst kompilieren musst.

Ralf9, 2019-01-04:
Diese hier ist für den nano cc1101
`set sduino flash https://raw.githubusercontent.com/Ralf9/SIGNALDuino/dev-r332_cc1101/firmware/SIGNALduino_nanoCC1101_3321rc6.hex`

HomeAutoUser, 2019-01-04:
Ich habe nun mal zurückgeschaut und konnte den Fehler wie bei @sidey79 nicht mit dem selben Bild erkennen. Auffällig waren dennoch diese Nachrichten welche immer wieder auftreten.

```
2019.01.04 21:56:28 4: sduino_433/msg READredu: MU;P0=-32001;P1=470;P2=-7831;P3=-1985;P4=-4009;CP=1;R=0;D=01212121212121212121212121212131314131314141314141313131414141314131313141314131413141314131413141313131314141414121213131413131414131414131313141414131413131314131413141314131413141314131313131414141412121313141313141413141413131314141413141313131413141;O;
2019.01.04 21:59:07 4: sduino_433/msg READredu: MU;P0=-32001;P1=472;P2=-7825;P3=-1980;P4=-3999;CP=1;R=3;D=01212121212121212121212121212131314131314141314141313131414141314131313141314131413141314131413141313131314141414121213131413131414131414131313141414131413131314131413141314131413141314131313131414141412121313141313141413141413131314141413141313131413141;O;
```

Ich vermute, das sind diese, welche bei @sidey79 den Fehler aufbringen und bei dir einen Overflow.

sidey79, 2019-01-04:
@HomeAutoUser 
Bitte mal diesen Build testen und die Meldungen wieder hier posten:

`set sduino_433 flash https://github.com/RFD-FHEM/SIGNALDuino/releases/download/nightly/SIGNALDuino_nanocc1101nightly.hex`

Grüße Sidey

HomeAutoUser, 2019-01-04:
@sidey79 wurde gemacht :-)

hier die spanischen Erkenntnisse
....

```
2019.01.05 00:17:40 4: sduino_433/msg READredu: MU;P0=-2004;P1=491;P2=-1001;P3=-4046;P4=-32001;D=1312121212101010101012121212121212101210101210101210101010121210101012121;CP=1;R=34;
2019.01.05 00:18:05 4: sduino_433/msg READredu: MU;P0=-2011;P1=490;P2=-4058;P3=-1000;P4=-32001;D=1210101010101010101013101313131313101013101313131010101010131310101313101;CP=1;R=24;
```
müsssten das nicht MS Nachrichten werden?

```
2019.01.05 00:18:51 4: sduino_433/msg READredu: MU;P0=-1954;P1=482;P2=-974;P3=-3904;D=01212121212101212121012101010101210121010101212131210121210121212101212101212121212121012121210121010101012101 00;CP=1;R=26;
2019.01.05 00:18:52 4: sduino_433/msg READredu: MU;P0=-4305;P1=386;P2=-2047;P3=-9632;D=0121210121210121012101212121212121212101212121012121010121312121 00;CP=1;R=220;
2019.01.05 00:19:28 4: sduino_433/msg READredu: MU;P0=-2078;P1=409;P3=-4617;P4=-9584;P5=-14912;D=010101313101410101310101310131013101010101 50;CP=1;R=220;P;
2019.01.05 00:19:28 4: sduino_433/msg READredu: MU;P0=-2078;P1=409;P3=-4617;P4=-9584;P5=-14912;D=010101313101410101310101310131013101010101 50;CP=1;R=220;P;
2019.01.05 00:19:41 4: sduino_433/msg READredu: MU;P0=-971;P1=493;P2=-1948;P3=836;P4=-3920;D=0121210101010101010101210121010121232121212101012121212121214101210101010101212101010101 00;CP=1;R=41;
2019.01.05 00:19:44 4: sduino_433/msg READredu: MU;P0=-11668;P1=476;P2=-3598;P3=-1959;D=01213121212121213131312131313131313131312121213121212121212131 20;CP=1;R=212;1;
2019.01.05 00:19:44 4: sduino_433/msg READredu: MU;P0=-380;P1=486;P2=-3920;P3=-1963;P4=-2564;D=0121212121213131312131313131313131312121213121212121 40;CP=1;R=213;
2019.01.05 00:19:45 4: sduino_433/msg READredu: MU;P0=-3909;P1=502;P2=-1989;D=0101010101212121012121212121212121010101210101010101 20;CP=1;R=212;
2019.01.05 00:19:45 4: sduino_433/msg READredu: MU;P0=-2780;P1=484;P2=-1847;P3=-3918;P4=-688;P5=-8548;D=0121313141513121313131313121212131212121212121212131 20;CP=1;R=211;1;
```

2 negative wieder hintereinander am Schluss

```
2019.01.05 00:19:25 4: sduino_433/msg READredu: MU;P0=-7356;P1=998;P2=-1027;P3=1338;P4=-635;P5=-466;P6=433;P7=-112;D=012341567121212121234121567121212341562651514651215621212651565;CP=1;R=224;V;V;
```
dort erstmals die Kürzel V;V gesehen sonst bei anderen Nachrichten nicht

sidey79, 2019-01-04:
Das Leerzeichen vor den doppelten, kamen die jetzt von dir?
Ich habe mir übrigens folgende Regex erstellt um die doppelten etwas schneller zu finden. 
https://regex101.com/r/o5fRVw/1/

Sowas wie `addValue` oder `after compress_pattern` hast Du nicht übergeben bekommen?

HomeAutoUser, 2019-01-04:
@sidey79 die Leerzeichen kamen von mir, sorry hatte ich nicht erwähnt. Sie sollten dir schneller zeigen wo die Stelle ist.

>Sowas wie addValue oder after compress_pattern hast Du nicht übergeben bekommen?

Sowas ist nicht im Logfile enthalten. Kein Übergabe!

sidey79, 2019-01-05:
Hmm, 

Ich  habe jetzt noch mal eine compiliert. Dort sollte CB; mit in der Zeile enthalten sein wenn es zu diesem Fehler kommt. Möchte prüfen, ob meine Prüfroutine funktioniert.

sidey79, 2019-01-05:
Flash / Download URL ist im übrigen identisch aber `compiled at Jan 5 2019 00:40:13`

HomeAutoUser, 2019-01-05:
Ich habe es soeben draufgeschoben.
Erste Erkenntnisse

```
2019.01.05 01:46:01 4: sduino_433/msg READredu: MU;P0=-2014;P1=482;P2=-1014;P3=-4060;P4=-32001;D=1312121212101010101012121212121212101210101212101010101010121210101010121;CP=1;R=34;
2019.01.05 01:46:21 4: sduino_433/msg READredu: MU;P0=-2004;P1=447;P3=-9216;P4=-3939;D=010131014101410141410101010141410101010101010100;CP=1;R=211;
2019.01.05 01:46:21 4: sduino_433/msg READredu: MU;P0=-1132;P1=445;P2=-2224;P3=-3962;D=0121213131212121212121212121312121313131213131212121312120;CP=1;R=211;
```

HomeAutoUser, 2019-01-05:
Ich bin dann erstmal im Fuchsbau ;-) G8

sidey79, 2019-01-05:
Hmm, das ist seltsam.

Mit dieser Routine prüfen ich den Puffer und es sollte ein CB ausgegeben werden wenn zwei Pulse mit gleichem Vorzeichen gefunden werden.
https://github.com/RFD-FHEM/SIGNALDuino/blob/c77158816713602d658086322a4908f7e7fe437e/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L46-L55

Offensichtlich klappt das ja nicht.

sidey79, 2019-01-05:
Ändert sich eigentlich etwas an der Situation wenn Du den MC oder MS decoder deaktivierst?

HomeAutoUser, 2019-01-05:
Moin Moin
>Ändert sich eigentlich etwas an der Situation wenn Du den MC oder MS decoder deaktivierst?

Ist dies bezogen auf die aktuell compilierte Version oder wenn ich den "Standard" nehme und meinst du das Verhalten mit doppelten negtiven?

HomeAutoUser, 2019-01-05:
nur MU aktiv!

doppter positiver
```
2019.01.05 12:03:29 4: sduino_433/msg READredu: MU;P0=491;P1=-1009;P2=804;P3=-2014;P4=-1332;P5=-120;P6=140;P7=-744;D=01232323040567010101030303030301010101010101030103030301010103030303010103030301030300;CP=0;R=31;

```

doppelter negativer
```
2019.01.05 12:03:48 4: sduino_433/msg READredu: MU;P0=-4055;P1=479;P2=-1006;P3=-2023;D=012121212131313131312121212121212131213131312121213131313121213131312131310121212121313131313121212121212121312131313121212131313131012121212131313131312121212121212131213131312121213131313121213131312130;CP=1;R=27;0;
2019.01.05 12:03:48 4: sduino_433/msg READredu: MU;P0=-32001;P1=479;P2=-1006;P3=-2023;P4=-4056;D=0121313131213131412121212131313131312121212121212131213131312121213131313121213131312130;CP=1;R=0;0;
2019.01.05 12:03:48 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2019.01.05 12:03:48 4: sduino_433/msg READredu: MU;P0=-9986;P1=394;P2=-2044;P3=-4622;P4=-3436;D=0101213131212131213121312121212121212131212121312121213131210121313121213121312131212121212121213121212140;CP=1;R=216;
2019.01.05 12:03:48 4: sduino_433/msg READredu: MU;P0=-2045;P1=401;P2=-4521;P3=-9620;D=012101310121210101210121012101010101010101210101012101010120;CP=1;R=216;
2019.01.05 12:03:48 4: sduino_433/msg READredu: MU;P0=-8960;P1=396;P2=-2042;P3=-4606;P4=-1360;D=0121313121213121312131212121212121213121212131212121313121012131312121312140;CP=1;R=219
```

sidey79, 2019-01-05:
Ich hab noch mal eine neue Firmware :)

HomeAutoUser, 2019-01-05:
ähm JA

 reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv reset nv re

sidey79, 2019-01-05:
So ich hab eine neue Firmware. Sollten jetzt wieder weniger aber dafür brauchbarere Informationen kommen.

HomeAutoUser, 2019-01-05:
Ich habe sie absofort laufen und sammel schon wieder.

```
---- zwei positive Pulse ----
2019.01.05 19:08:29 4: sduino_433/msg READredu: MU;P0=850;P1=533;P2=-25136;P3=1340;P4=-1116;P5=-837;P6=-5864;D=123434343414341534343434341414143414343414341434143414141434341414341434143414141414143436123434343414350434343434341414143414343414341434143504141434341414341434143414141414143430;CP=1;R=214;0;4;0;

---- zwei negative Pulse ----
2019.01.05 19:09:08 4: sduino_433/msg READredu: MU;P0=-368;P1=133;P2=-113;D=012121212121212121212121212121212121212120;CP=1;R=236;0;
```

... ich ergänze hier weiter

HomeAutoUser, 2019-01-05:
Ich weiß nicht ob es zum Ansatz reicht oder was du gezielt haben möchtest.
Anbei mal das komplette Log bis zu dem jetzigen Zeitpunkt wo ich die Daten entnommen habe.
[Bug_searching.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/2729785/Bug_searching.txt)

sidey79, 2019-01-05:
Kannst Du auch noch einmal ohne die Message Reduktion so ein Log anfertigen?

Ich weiss selbst noch nicht so genau wo der Fehler steckt um die Laufzeit nicht zu sehr zu verändern habe ich halt nur wenig Debug Meldungen eingebaut.
Seltsam ist, dass die Überprüfung des Puffers nie einen Fehler findet.

HomeAutoUser, 2019-01-05:
Ich habe nun mal die Message Reduktion ausgeschalten vor fast 25minuten und habe den Eindruck hier ist es weniger. Dies kann aber täuschen weil ich so viel immer scrollen muss. Es ist aber auch noch vorhanden mit doppelten gleichen Pulsen.

```
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 82 -> Fernotron matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868/msg READredu: MU;P0=-141;P1=750;P2=-305;P3=190;P5=358;P6=533;P7=-620;D=0123012305062101030306067523212306060323010500;CP=3;R=212;
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 19 -> minify matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 73 -> FHT80 matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 80 -> EM1000WZ matches, trying to demodulate
2019.01.05 20:18:41 4: sduino_868: Fingerprint for MU Protocol id 82 -> Fernotron matches, trying to demodulate
2019.01.05 20:18:42 4: sduino_868/msg READredu: MU;P0=-143;P1=198;P2=357;P3=-800;P4=753;P5=-296;P7=-460;D=010102023451045202040201015204010171020101545104510200;CP=1;R=211;
2019.01.05 20:18:42 4: sduino_868: Fingerprint for MU Protocol id 19 -> minify matches, trying to demodulate
```
Anbei das Logfile [Uploading Bug_searching_without_CDR.txt…]()

sidey79, 2019-01-05:
@HomeAutoUser 
Hast Du den sduino_868 auch geflasht oder nur den 434?

Ich finde nur Debug Meldungen von dem 434, dafür aber keine doppelten gleichen Pulse.
Auf doppelte Vorzeichen habe ich noch nicht vollständig geprüft.

HomeAutoUser, 2019-01-05:
@sidey79 ich habe vorerst nur den 433 geflasht. Den anderen kann ich gern auch mal dort zum testen flashen.

>Auf doppelte Vorzeichen habe ich noch nicht vollständig geprüft.

Also ist das wohl die erste Erkenntnis, das im Code noch nicht auf sowas geprüft wurde?

sidey79, 2019-01-05:
> @sidey79 ich habe vorerst nur den 433 geflasht. Den anderen kann ich gern auch mal dort zum testen flashen.

Muss nicht sein. Ich wollte nur sichergehen.

> > Auf doppelte Vorzeichen habe ich noch nicht vollständig geprüft.
> Also ist das wohl die erste Erkenntnis, das im Code noch nicht auf sowas geprüft wurde?

Doch eigentlich schon.  Ich suche aktuell mit meinen Augen :)
Findest Du im Logfile "Bug_searching_without_CDR.txt" eine fehlerhafte Übertragung für den sduino_434? Ich habe noch nichts gefunden.

Was ich suche sind MU Ausgaben die einen Fehler enthalten und in der Nächsten empfangenen Zeile steht dann hoffentlich eine Debug Meldung nv reset

Hast Du zu dieser Meldung noch mehr Daten?
`2019.01.05 19:09:54 4: sduino_433/msg READredu: MU;P0=498;P1=-1000;P2=-2013;P3=-4000;P4=96;P5=-94;D=010101020302010101454202020301010202454540;CP=0;R=235;0;E;
`

HomeAutoUser, 2019-01-05:
> Hast Du zu dieser Meldung noch mehr Daten?
> `2019.01.05 19:09:54 4: sduino_433/msg READredu: MU;P0=498;P1=-1000;P2=-2013;P3=-4000;P4=96;P5=-94;D=010101020302010101454202020301010202454540;CP=0;R=235;0;E; `

Bitteschön.

```
2019.01.05 19:09:39 4: sduino_433: Fingerprint for MU Protocol id 80 -> EM1000WZ matches, trying to demodulate
2019.01.05 19:09:39 4: sduino_433: Fingerprint for MU Protocol id 82 -> Fernotron matches, trying to demodulate
2019.01.05 19:09:39 4: sduino_433/msg READredu: MS;P0=-1836;P1=522;P2=-1014;P3=750;P4=-4024;P5=-774;D=141012101210101012101212121212153212101212101212101010101012101210101010;CP=1;SP=4;R=217;
2019.01.05 19:09:39 4: sduino_433: Matched MS Protocol id 7 -> weather
2019.01.05 19:09:54 4: sduino_433/msg READ:  nv reset-116,-2052,32,32
2019.01.05 19:09:54 4: sduino_433/msg READ: 
2019.01.05 19:09:54 4: sduino_433/msg READ: Sync: -4704 -> SyncFact: -9.82, Clock: 479, Tol: 415, PattLen: 8 (0), Pulse: -116, -2052, mStart: 0, MCD: 0, mtrunc: 1
2019.01.05 19:09:54 4: sduino_433/msg READ: Signal: 04030403030303020203030403030202.  [32]
2019.01.05 19:09:54 4: sduino_433/msg READ: Pattern:  P0: 4*[479] P1: 0*[-1468] P2: 2*[-2052] P3: 1*[-1003] P4: 1*[-4704] P5: 3*[100] P6: 2*[-94] P7: 0*[]
2019.01.05 19:09:54 4: sduino_433/msg READredu: MU;P0=498;P1=-1000;P2=-2013;P3=-4000;P4=96;P5=-94;D=010101020302010101454202020301010202454540;CP=0;R=235;0;E;
2019.01.05 19:09:54 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
2019.01.05 19:09:54 4: sduino_433: Fingerprint for MU Protocol id 61 -> FS10 matches, trying to demodulate
2019.01.05 19:09:54 4: sduino_433: Fingerprint for MU Protocol id 70 -> FHT80TF matches, trying to demodulate
2019.01.05 19:09:54 4: sduino_433: Fingerprint for MU Protocol id 74 -> FS20 matches, trying to demodulate
2019.01.05 19:09:54 4: sduino_433: Fingerprint for MU Protocol id 80 -> EM1000WZ matches, trying to demodulate
2019.01.05 19:09:55 4: sduino_433/msg READredu: MU;P0=100;P1=-2112;P2=498;P3=-1016;P6=-116;P7=-4508;D=121232321212123230601232323232723212323;CP=2;R=237;0;2;0;
2019.01.05 19:09:55 4: sduino_433: Fingerprint for MU Protocol id 8 -> TX3 Protocol matches, trying to demodulate
```

>Findest Du im Logfile "Bug_searching_without_CDR.txt" eine fehlerhafte Übertragung für den sduino_434? Ich habe noch nichts gefunden.

Ich schaue nochmal rein.

sidey79, 2019-01-05:
Ich bin nicht sicher, ob wir nicht bei READredu auch noch einen Bug haben.
Lass uns doch erst mal ohne Message Reduction suchen.

Wenn wir uns der Message Reduction annehmen, dann sollten wir auch die übergebenen Meldungen auf der FHEM Seite loggen, Aktuell passiert das leider nur, wenn debug=1.

HomeAutoUser, 2019-01-06:
Ich habe nun die Reduction eingeschalten und sofort erhalte ich auf dem Empfänger wieder Fehler mit gleichen Pulsen.

```
ERROR - same Pulse 1
2019.01.06 00:55:35 4: sduino_868/msg READredu: MU;P0=-122;P1=98;P3=-207;P4=203;P5=-94;D=01010131343434343401010101315401010134015100;CP=1;R=214;
ERROR - same Pulse 2
2019.01.06 01:38:49 4: sduino_433/msg READredu: MU;P0=-1000;P1=495;P2=-2017;P3=-4060;D=010121210101212101012121212101012121012121213121212121212121212101210101010101212100;CP=1;R=0;0;
ERROR - same Pulse 3
2019.01.06 01:39:09 4: sduino_433/msg READredu: MU;P0=-1009;P1=493;P2=-2007;P3=840;P4=-644;P5=-4080;D=0101010101212101010101212121210123412101210121512101210121212101210101010101010100;CP=1;R=218;0;
ERROR - same Pulse 4
2019.01.06 01:39:10 4: sduino_433/msg READredu: MU;P0=489;P1=-2007;P2=-1003;P3=-4060;P4=824;P6=-468;P7=-92;D=01020202020101010102010201020102010341020102010101020102020202020202020201010206070202010101010201020102010200;CP=0;R=217;0;
ERROR - same Pulse 5
2019.01.06 01:39:11 4: sduino_433/msg READredu: MU;P0=735;P1=-424;P2=350;P3=-797;P4=220;P5=-284;P6=-156;D=01234521010103210101032323210101010321032321010101010321010103260321010103210300;CP=0;R=236;0;
ERROR - same Pulse 6
2019.01.06 01:39:11 4: sduino_433/msg READredu: MU;P0=735;P1=-424;P2=-797;P3=351;P4=-4008;P5=1000;P6=-156;D=01023101010232310431010102310101023232310101010231023231010101560231010102310200;CP=0;R=236;0;
ERROR - same Pulse 7
2019.01.06 01:39:12 4: sduino_433/msg READredu: MU;P0=407;P1=-4414;P2=-102;P3=-1976;P5=605;P7=-628;D=010203020303535353530703035153530103515353530300;CP=0;R=222;0;
ERROR - same Pulse 8
2019.01.06 01:39:12 4: sduino_433/msg READredu: MU;P0=-254;P1=-2098;P2=387;P3=-1344;P4=-500;P5=-3872;P7=-388;D=121212324212321242324252421232721232025200;CP=2;R=222;0;R;
ERROR - same Pulse 9
2019.01.06 01:39:24 4: sduino_433/msg READredu: MU;P0=592;P1=-1524;P3=-2173;P4=-4201;P5=448;P6=-244;P7=-9120;D=010303030303040304030304015603030304030403030700;CP=0;R=221;0;
ERROR - same Pulse 10
2019.01.06 01:39:25 4: sduino_433/msg READredu: MU;P0=-2141;P1=587;P2=-4187;P3=-885;P4=448;P5=1034;P7=-6652;D=01010101010101010121012134310121010101010535012101010471210101210101012101040121010101010100;CP=1;R=0;0;
ERROR - same Pulse 11
2019.01.06 01:39:26 4: sduino_433/msg READredu: MU;P0=240;P1=-2080;P2=359;P3=-3943;P4=572;P5=-9276;P6=744;D=0123232123212323412125032321212123212163212321212100;CP=2;R=210;0;
ERROR - same Pulse 12
2019.01.06 01:39:27 4: sduino_433/msg READredu: MU;P0=266;P1=-3980;P2=424;P3=-1995;P4=-528;P5=828;P6=-8240;D=0123212324562123212121212123232321232323232323232123212323212121232321232123242121212300;CP=2;R=211;0;
ERROR - same Pulse 13
2019.01.06 01:39:32 4: sduino_433/msg READredu: MU;P0=-2019;P1=494;P2=-999;P3=-4060;D=0121012101010101012101010101212101010101012131212121210101010101212121212121210100;CP=1;R=26;0;
ERROR - same Pulse 14
2019.01.06 01:39:33 4: sduino_433/msg READredu: MU;P0=-2010;P1=496;P2=-986;P3=-4060;D=0121010101010121212121212121012101010101012101010101212101010101012131212121210100;CP=1;R=23;0;
ERROR - same Pulse 15
2019.01.06 01:39:37 4: sduino_433/msg READredu: MU;P0=-2005;P1=481;P2=-1013;P3=-4048;D=012121212101012101212101010101010121210101012101213121010121212121010121210121212100;CP=1;R=20;0;
ERROR - same Pulse 16
2019.01.06 01:39:38 4: sduino_433/msg READredu: MU;P0=-2004;P1=508;P2=-991;P3=-4044;D=012121212101012121012121212101012101212101010101010121210101012101213121010121212121010121210121212121010121012121010101010101212101012121212101012100;CP=1;R=25;0;
ERROR - same Pulse 17
2019.01.06 01:39:39 4: sduino_433/msg READredu: MU;P0=375;P1=-2093;P3=-4620;P4=848;P5=-9648;P7=-1576;D=0103030101030101030103030301010101010103010103430301030101050301070101030101030100;CP=0;R=209;0;
ERROR - same Pulse 18
2019.01.06 01:39:41 4: sduino_433/msg READredu: MS;P0=-976;P2=-3892;P3=486;P4=-1944;D=323034303030303034343030303030303030343430303430343434343430343030303034303434303030300;CP=3;SP=2;R=43;
ERROR - same Pulse 19
2019.01.06 01:39:47 4: sduino_433/msg READredu: MU;P0=-969;P1=487;P2=-1942;P3=-3892;D=0101210101010101012101210101012121212101210121012121013101210101210101012101012100;CP=1;R=36;0;
ERROR - same Pulse 20
2019.01.06 01:39:48 4: sduino_433/msg READredu: MU;P0=387;P1=-4663;P2=-2319;P3=520;P4=-7376;P5=-700;P6=1034;P7=-942;D=010201020202020202020201023201010102010102040202010101020102056767020200;CP=0;R=209;0;
ERROR - same Pulse 21
2019.01.06 01:40:06 4: sduino_433/msg READredu: MS;P0=-1022;P1=-312;P2=-4066;P3=479;P4=-1863;P5=745;D=32343034305434343034303030303030303030343430303030343031343430343034303430343430345030300;CP=3;SP=2;R=224;
ERROR - same Pulse 22
2019.01.06 01:40:06 4: sduino_433/msg READredu: MU;P0=-2012;P1=495;P2=-1005;P3=-3964;D=01210121013101210121010101210121212121212121212101012121212101010101210121012100;CP=1;R=218;0;
ERROR - same Pulse 23
2019.01.06 01:40:06 4: sduino_433/msg READredu: MU;P0=-1833;P1=559;P2=-991;P3=-404;P4=272;P5=-304;D=0121010101210121212121345121212121010121212121010100;CP=1;R=220;0;
ERROR - same Pulse 24
2019.01.06 01:40:06 4: sduino_433/msg READredu: MU;P0=-1560;P1=1012;P2=-1006;P3=493;P4=-2010;P5=-146;P6=-4056;D=01232323232323232323434323232323434343530323432343234323436343234323434343235303232323232323232323434323232323434343432300;CP=3;R=218;0;
ERROR - same Pulse 25
[... cut ...]
ERROR - same Pulse 37
2019.01.06 01:41:05 4: sduino_433/msg READredu: MU;P0=134;P1=-3994;P2=381;P3=-2058;P4=628;P5=-9244;P6=288;D=012321232321212321232141232325612123232300;CP=2;R=0;0;
ERROR - same Pulse 38
2019.01.06 01:41:05 4: sduino_433/msg READredu: MU;P0=-1578;P1=397;P2=-7920;P3=-904;P4=-4009;P6=-2052;D=0141614161414161612131414161616141616141614100;CP=1;R=211;0;A;

```

@sidey79 wir müssen die Reduction unter die Lupe nehmen wo da die Fehler auftreten.
Mit dieser Erkenntnis ist Schluss für heute.

sidey79, 2019-01-06:
Habe bezüglich der Message Reduction etwas angepasst und eine neue FW erstellt.
Müssen wir uns halt noch ein wenig weiter vortasten :)

HomeAutoUser, 2019-01-06:
Ich war mal so fix vor dem Ausschalten :-)
Hier die Fehler von wenigen Minuten mit deiner neuen FW und das Logfile dazu. [Bug_searching_new_test.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/2730031/Bug_searching_new_test.txt)

```
ERROR same Pulse 1
2019.01.06 01:54:09 4: sduino_433/msg READredu: MU;P0=426;P1=-2023;P2=-4604;P3=-436;P4=-9616;P5=-280;P6=652;P7=-2684;D=03010104020102010202050201020201020101010101016201020102010102010102020201020102020201010102010201010101010102010700;CP=0;R=0;
ERROR same Pulse 2
2019.01.06 01:54:09 4: sduino_433/msg READredu: MU;P0=-992;P1=505;P2=-2009;P3=-4072;D=0101210101010121210121010121212121212101012121210121013101212101010101212101012101010100;CP=1;R=0;
ERROR same Pulse 3
2019.01.06 01:54:17 4: sduino_433/msg READredu: MU;P0=166;P1=-2086;P2=351;P3=-4026;P4=672;P6=-8060;P7=-732;D=012121232123212143232123212323212126272323212121232100;CP=2;R=0;
ERROR same Pulse 4
2019.01.06 01:54:21 4: sduino_433/msg READredu: MU;P0=484;P1=-989;P2=648;P3=-2015;P4=-452;D=0101010101210101010303010101010303030301030400;CP=0;R=0;
ERROR same Pulse 5
2019.01.06 01:54:21 4: sduino_433/msg READredu: MU;P0=-596;P1=495;P2=-1000;P3=-2025;P4=656;P5=-4064;D=01212121313131312131213121312134513121312131313121312121212100;CP=1;R=0;
ERROR same Pulse 6
2019.01.06 01:54:50 4: sduino_433/msg READredu: MU;P0=227;P1=-2067;P2=367;P3=-1104;P4=-536;P5=-3953;P6=-112;P7=-9268;D=01212121232425212521212525212521252625212127052521212100;CP=2;R=0;
ERROR same Pulse 7
2019.01.06 01:54:50 4: sduino_433/msg READredu: MU;P0=-9192;P1=-4056;P2=422;P3=-803;P4=-2054;P5=-2880;P7=1048;D=12323242124212424242424242421242325242421212421242123742424200;CP=2;R=0;B;
ERROR same Pulse 8
2019.01.06 01:54:51 4: sduino_433/msg READredu: MU;P0=122;P1=-3986;P2=430;P3=-2048;P4=788;P5=-1216;D=0121232323412323232323232321232123232325212123232100;CP=2;R=0;
```

So kannst du mit dem Zeitstempel genau die Zeile wiederfinden mit allen Informationen davor und dahinter.

Gute Nacht

sidey79, 2019-01-06:
N8.

In Zeile 1388 oder so in etwa von der 00_SIGNALduino.PM

Die müsste geändert werden zum debuggen:
`  SIGNALduino_Log3 $name, 4, "$name/RAW READ: $SIGNALduinodata/$buf";  `

sidey79, 2019-01-06:
Nach dem Sichten von Logfiles bin ich der Meinung, dass der Fehler im FHEM Modul steckt.
Wo weiss ich leider noch nicht. Aber immer wenn eine gerade Anzahl übergeben wird, dann wird ein Byte zu viel ausgewertet

Ralf9, 2019-01-06:
Hier das D und d tauschen müsste reichen, siehe auch mein vorheriger comment

https://github.com/RFD-FHEM/SIGNALDuino/blob/22d94f7bbe3fe3ab2e3df5cafef203dac7c565a4/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L718-L723

sidey79, 2019-01-06:
Der Datenteil selbst wird von dem Microcontroller ohne die doppelten Indexe übergeben.
Die kommen erst bei der Umwandlung im Modul zustande.

sidey79, 2019-01-06:
Ich erinnere mich, dass es zu dieser Verwechslung schon einmal eine Diskussion gab.
Einen Commit dazu habe ich auf die schnelle nicht gefunden.

Es erscheint mir auch besser die Anpassungen auf der Fhem Seite vorzunehmen.
Dann passt es zu bereits installierten Firmwares

sidey79, 2019-01-06:
@HomeAutoUser 

Damit sollten die doppelten nullen am Ende nicht mehr auftauchen.

HomeAutoUser, 2019-01-06:
@Ralf9, welche Nebenwirkungen und wenn du dies äußerst , Klär uns bitte auf wenn wir ggf vorher etwas verhindern können.

Ralf9, 2019-01-06:
Solch eine versuchte Fehlerbeseitigung im fhem Modul möchte ich nicht unterstützen.
Der bug ist in der firmware und da sollte er auch behoben werden.
Wenn das d und D hier getauscht wird, hat es keine Nebenwirkungen
https://github.com/RFD-FHEM/SIGNALDuino/blob/22d94f7bbe3fe3ab2e3df5cafef203dac7c565a4/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L718-L723

HomeAutoUser, 2019-01-06:
@sidey79 ich muss in der Beziehung @Ralf9 zustimmen. Einen Fehler sollte man beseitigen an der Stelle wo er Auftritt und nicht etwas anderes machen um diesen auszugleichen. Der Wille zu einer fehlerfreien Firmware ist unser Ziel, lass uns es bitte ergründen auch wenn es schwierig ist. Vielleicht findet man auch dadurch noch andere Kleinigkeiten.

sidey79, 2019-01-06:
@HomeAutoUser  @Ralf9 
Ich verstehe nicht, wieso der Fehler mit den doppelten identischen Pulsen am Ende einer Nachricht eurer Meinung in der Firmware liegt und nicht im Modul.

In die Firmware wurde vor langer Zeit der Code eingebaut
Für MU Nachrichten:
https://github.com/RFD-FHEM/SIGNALDuino/commit/d212fbff21d024cca03a05dbb9ddbc4e64741f9f
Für MS Nachrichten:
https://github.com/RFD-FHEM/SIGNALDuino/commit/8216f67ff6639542e571fe37cfc5f7f4dee686d8

In beiden Fällen wurde folgendes programmiert:

```
					if ((messageLen & 1) == 1) {   // ungerade
						MSG_PRINT("D");
					}
```
'D' wird also für ungerade und 'd' für eine gerade Anzahl an Bits verwendet.

**Sämtliche Firmware, die seit dem compiliert wurde (2 Jahre) verwendet also diese Festlegung.**

Im FHEM Modul wurde die Implementierung dazu nicht passend vorgenommen: :(
https://github.com/RFD-FHEM/RFFHEM/commit/7210cc0d4b85eb8b12f3b22664e0ea6e278012ca

Hier wurde angenommen, bei 'd' wird eine ungerade Anzahl an Bits gesendet:
```
				if ($m0 eq "d") {
					$partD =~ s/.$//;	   # letzte Ziffer entfernen wenn Anzahl der Ziffern ungerade
				}
```

Ob damals nun d oder D für eine ungerade Anzahl die beabsichtige Variante war lässt sich nicht mehr feststellen.
Die Anpassung des FHEM Moduls auf die Variante 'D' für ungerade wäre sowohl in der SVN Version als auch in der DEV-R33 Version problemlos machbar, da sämtliche Firmware die schon compiliert wurde damit ordnungsgemäß funktionieren würde.

Ralf9, 2019-01-06:
Das Problem ist aber, daß der bug nur bei den MU-Nachrichten ist,
wenn Du d und D im fhem Modul tauscht, dann passt es bei den MS-Nachrichten nicht mehr.

sidey79, 2019-01-06:
Hmm sowohl im MS als auch im MU Code wird D für ungerade verwendet.
Was meinst Du?

HomeAutoUser, 2019-01-06:
@sidey79 die schwere des Bugs kann ich nicht so einschätzen wie @Ralf9.
Meine Meinung war auf jedenfall neutral zu betrachten und das ich immer für einen Ansatz der Behebung bin wo dies auch passiert und nicht zum ausbügeln an anderer Stelle dies richtet.

Das der Fehler nur bei MU Nachrichten vorkommt ist FALSCH!

Ich habe vom Heutigen Tag bis jetzt geprüft! Folgendes kam heraus:

- read 444343 lines
- found 10208 error
- MU 10170
- MS 38

```
ERROR same Pulse 165
2019.01.06 02:08:49 4: sduino_433/msg READredu: MS;P0=-1940;P1=503;P6=-3753;P7=-964;D=16171017171017171710171710171717171717101717101010101010101710171017101717171017171017177;CP=1;SP=6;R=34;
....
ERROR same Pulse 166
2019.01.06 02:09:06 4: sduino_433/msg READredu: MU;P0=-2010;P1=499;P2=-1004;P3=-4064;D=0121010101010121210101210101213101010101010101010121012121212121010121210100;CP=1;R=0;
....
ERROR same Pulse 224
2019.01.06 02:14:50 4: sduino_433/msg READredu: MS;P0=-969;P2=-3886;P3=493;P4=-1957;D=32303430303030303434303030303030303034343030343034343434343034303030303030343430303030303030300;CP=3;SP=2;R=45;
ERROR same Pulse 225
2019.01.06 02:14:54 4: sduino_433/msg READredu: MU;P0=296;P1=-2045;P2=536;P3=-4205;P4=1004;P6=-6808;D=012321212343232121212123212321212126232100;CP=2;R=0;
```

Ich kann auch gegentesten mit der jetzigen Anpassung aber wie schon oben gesatzt bitte Behebung ja aber an der richtigen Stelle.

Ralf9, 2019-01-06:
Ich habe es mir nochmals angeschaut.
- Bei MS zeigt `mend` auf den letzten Wert der Nachricht z.B.
D=1234 und in der message[] sind es die Positionen `0123` also mend ist mit 3 ungerade also "D"
- Bei MU zeigt messageLen hinter das letzte Zeichen
es hätte also heissen müssen:
`if ((messageLen-1 & 1) == 1) { // ungerade`

[27 more replies]

closed 2020-01-13 by HomeAutoUser