RFD-FHEM/SIGNALDuino #107: dev-r33_cc1101-espmerge - to big for radino

elektron-bbs, 2019-01-11:
Die Firmware ist jetzt zu groß für den Radino geworden. Lässt sich da noch etwas optimieren?

Der Sketch verwendet 29100 Bytes (101%) des Programmspeicherplatzes. Das Maximum sind 28672 Bytes.
Globale Variablen verwenden 1040 Bytes des dynamischen Speichers.

sidey79, 2019-01-11:
welche Fehlermeldung kommt denn mit dem richrigen Board?

elektron-bbs, 2019-01-12:
Ich habe jetzt nochmal die vorhergehende (mit kleinen Änderungen von mir) mit der jetzigen verglichen. Die aktuelle bracht knapp 3 kB mehr Flash. Für den Radino brauchte es immer so etwa 2 kB mehr als beim Nano (wahrscheinlich für den USB).  Das kann dann bei der aktuellen Version nicht mehr funktionieren. Vieleicht liegt es ja tatsächlich nur an der Größe.

nanoCC1101 - SIGNALDuino-dev-r33_cc1101_2018-11-19_unused-pins
Der Sketch verwendet 25450 Bytes (82%) des Programmspeicherplatzes. Das Maximum sind 30720 Bytes.
Globale Variablen verwenden 1027 Bytes (50%) des dynamischen Speichers, 1021 Bytes für lokale Variablen verbleiben. Das Maximum sind 2048 Bytes.

radinoCC1101 - SIGNALDuino-dev-r33_cc1101_2018-11-19_unused-pins
Der Sketch verwendet 27868 Bytes (97%) des Programmspeicherplatzes. Das Maximum sind 28672 Bytes.
Globale Variablen verwenden 1008 Bytes des dynamischen Speichers.

nanoCC1101 - SIGNALDuino-dev-r33_cc1101-espmerge_2019-01-11
Der Sketch verwendet 28146 Bytes (91%) des Programmspeicherplatzes. Das Maximum sind 30720 Bytes.
Globale Variablen verwenden 1259 Bytes (61%) des dynamischen Speichers, 789 Bytes für lokale Variablen verbleiben. Das Maximum sind 2048 Bytes.

elektron-bbs, 2019-01-13:
Die Ausgabe des verwendeten Chips bei "get Version" könnte man nur beim ESP compilieren lassen - spart etwa 150 Byte:

commands.h
#ifdef PIN_MARK433
        MSG_PRINT("(");
				MSG_PRINT(FPSTR(isLow(PIN_MARK433) ? TXT_433 : TXT_868));
				MSG_PRINT(FPSTR(TXT_MHZ));
        MSG_PRINT(")");
#else
	#ifdef CMP_CC1101
    #ifdef ESP8266
        MSG_PRINT("("); MSG_PRINT(FPSTR(TXT_CHIP)); MSG_PRINT(FPSTR(TXT_BLANK)); MSG_PRINT(FPSTR(TXT_CC110));
				switch (cc1101::chipVersion()) {
					case 0x03:
						 MSG_PRINT("0"); 
						break;
					case 0x14:
					case 0x04:
						 MSG_PRINT("1");
						break;
					case 0x05:
						 MSG_PRINT("0E");
						break;
					case 0x07:
						 MSG_PRINT("L");
						break;
						break;
					default:
						MSG_PRINT(" unknown");
						break;
				}
				MSG_PRINT(")");
    #endif
	#endif
#endif

Für den Radino reicht das aber immer noch nicht :-(

sidey79, 2019-01-13:
@elektron-bbs 

Ich habe alles neu compiliert. Reicht nun auch für den Radino, der scheinbar mehr braucht als der pro micro:

```
Sketch uses 28490 bytes (99%) of program storage space. Maximum is 28672 bytes.
Global variables use 962 bytes of dynamic memory.
Compile radino cc1101 finshed
```

elektron-bbs, 2019-01-14:
Reicht aber gerade so :-) Hab ich jetzt geflasht und fnktioniert.
Es gibt aber scheinbar noch ein weiteres Problem. Ich mache ein neues Issue auf.

closed 2019-01-14 by sidey79