RFD-FHEM/SIGNALDuino #108: SIGNALDuino-dev-r33_cc1101-espmerge_2019-01-14 - schlechter Empfang

elektron-bbs, 2019-01-14:
Ich habe jetzt aktuell hier 4 Stück SIGNALduino mit dieser Firmware laufen. Bei 3 Stück auf 434 Mhz eigentlich keine Auffälligkeiten. Bei einem Nano auf 868 MHz funktioniert der Empfang der FHT-Temperaturregler damit nur noch unzureichend. In ca. 2,5 h wurden nur 7 Pakete dekodiert. Der Regler sendet eigentlich etwa alle 2 Minuten.
Nach Einspielen der älteren Version
V 3.3.1-RC-X-171218 SIGNALduino cc1101  - compiled at Dec 18 2018 20:44:14
ohne irgendwelche anderweitigen Änderungen wird der Regler wieder normal empfangen.
Mir kommt es so vor, als ob der SIGNALduino jetzt mit den etwas längeren Protokollen (knapp 70 Bit) nicht klar kommt. Die MU-Daten sind fast immer zu kurz. Anbei das Log:
[fhem_2019-01-14.log](https://github.com/RFD-FHEM/SIGNALDuino/files/2756816/fhem_2019-01-14.log)

sidey79, 2019-01-14:
Ich hab mal das Diff zwischen Jetzt und dem 18.12 erstellt:
https://github.com/RFD-FHEM/SIGNALDuino/compare/8bcd681d2e68bc4001646049c1638a20676be1b4...dev-r33_cc1101-espmerge

sidey79, 2019-01-14:
@elektron-bbs 

Es gibt schon ein paar Korrekturen, sieht man auch im diff. Relevant könnten die Register Einstellungen sein.
Seit dem 19. Dezember sind die von dir angepassten Werte enthalten. Kann es damit zusammenhängen?

elektron-bbs, 2019-01-15:
Nachtrag:
Nach genauerer Analyse der Logs habe ich jetzt auch festgestellt, das nicht nur der RSSI fehlt, sondern in diesen Zeiten auch keine anderen Sensordaten dekodiert werden. Es betrifft auch Sensoren, die sehr gute Empfangsbedingungen haben.

sidey79, 2019-01-15:
Hättest Du zwischendurch andere nightly Versionen Im Einsatz dir den Fehler nicht hätten?
Ich würde versuchen den Zeitraum des Fehlers einzuschränken.
Also ggf Mal eine Zwischenversion compilieren und dann weiter schauen.

elektron-bbs, 2019-01-15:
Angefangen hatte ich mit den Versuchen mit
2019.01.11 17:14:01 3: sduino434: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-10.hex provided, trying to flash
Damit fingen die Probleme an.
Ich werde jetzt mal Stück für Stück rückwärts probieren.

sidey79, 2019-01-15:
Wenn Du selber compiliert bekommst erleichtert das natürlich etwas die Fehlersuche.

Ich persönlich würde halt immer mal ein paar commits zurück springen. Dann sollte sich das Problem leichter einkreisen können

elektron-bbs, 2019-01-15:
Kann ich die Nightly-Builds auch irgendwo herunter laden?
Ich bin jetzt erstmal tageweise zurück gegangen und habe selbst compiliert:
2019.01.15 20:55:35 3: sduino868: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-06.hex provided, trying to flash
schlechter Empfang FHT
2019.01.15 21:18:20 3: sduino868: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-05.hex provided, trying to flash
kein Empfang FHT
2019.01.15 21:28:18 3: sduino868: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-04.hex provided, trying to flash
schlechter Empfang FHT

sidey79, 2019-01-15:
aktuell kannst Du die nicht herunter laden, ich kann es aber compilieren "lassen"

sidey79, 2019-01-15:
Ich hab mal einen etwas älteren Stand compilieren lassen:
https://github.com/RFD-FHEM/SIGNALDuino/releases/edit/nightly-20181219

elektron-bbs, 2019-01-16:
Ich glaube, so weit zurück gehen müssen wir vieleicht nicht.
Vorhin habe ich den Stand vom 03.01.2019 compiliert und geflasht. Damit funktioniert der Empfang des FHT noch, während mit Stand vom 04.01.2019 fast nichts ging. Ich sehe zwar gerade dort keine relevanten Änderungen, aber da blickst du besser durch. Der Test läuft jetzt allerdings erst seit gut 2,5 h. Morgen weiß ich mehr.

2019.01.15 20:55:35 3: sduino868: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-06.hex provided, trying to flash
schlechter Empfang FHT
2019.01.15 21:18:20 3: sduino868: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-05.hex provided, trying to flash
kein Empfang FHT
2019.01.15 21:28:18 3: sduino868: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-04.hex provided, trying to flash
schlechter Empfang FHT
2019.01.16 17:51:25 3: sduino868: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-03.hex provided, trying to flash
seit ca. 2,5 h Empfang normal
2019.01.16 17:58:42 3: sduino434: filename ./FHEM/firmware/SIGNALDuino-dev-r33_nanocc1101-espmerge_2019-01-03.hex provided, trying to flash
seit ca. 2,5 h Empfang normal

sidey79, 2019-01-16:
@elektron-bbs 
Welcher Commit?

elektron-bbs, 2019-01-16:
Der funktionierende:
https://github.com/RFD-FHEM/SIGNALDuino/commit/c6ef9287f85f01f74a4981aa855839f0a4d2f375
Der nicht mehr funktionierende:
https://github.com/RFD-FHEM/SIGNALDuino/commit/3fc208b46bfe23195c5af7f2b6ec01d4b86acdc2

sidey79, 2019-01-16:
Mich überrascht, dass diese Änderungen relevant sein sollen. Wäre Interessant ob es hilft, die checkMBuffer Abfrage in dem aktuelleren commit zu deaktivieren

elektron-bbs, 2019-01-17:
Vorab zur Info: Nach etwa 24 h gab es keine Aussetzer mit der Version vom 03.01.2019 bei beiden SIGNALduinos. 

Ich habe jetzt in der Version vom 04.01.2019 in der 
void SignalDetectorClass::addData(const int8_t value)
dieses auskommentiert:

	if (message.addValue(value))
	{
		messageLen++;
		// if (!checkMBuffer())
		// {
			// SDC_PRINT(" addValue ->");
			// SDC_PRINT("val="); SDC_WRITE(value);
			// SDC_PRINT(" msglen="); SDC_WRITE(messageLen);
			// SDC_PRINT(" bytc="); SDC_WRITE(message.bytecount);
			// SDC_PRINT(" valc="); SDC_WRITE(message.valcount);
			// SDC_PRINT(" mTrunc="); SDC_WRITE(m_truncated);
			// SDC_PRINT(" state="); SDC_WRITE(state);
			// SDC_PRINT(" success="); SDC_WRITE(success);
			// SDC_PRINT(" fir="); SDC_WRITE(*first);
			// SDC_PRINT(" las="); SDC_WRITE(*last);
			// SDC_PRINTLN("");
		// }
		m_truncated = false; // Clear truncated flag

	}

Test läuft jetzt.

sidey79, 2019-01-17:
Ich bin gespannt. Könnte aber in der Tat ein Problem mit der Verarbeitungszeit sein.

elektron-bbs, 2019-01-17:
Es sieht ganz danach aus.
In den vergangenen 3 h war die Dekodierung einwandfrei. Danach habe ich nochmal eine Gegenprobe gemacht und die Zeilen wieder aktiviert, es wurden nur noch wenige Nachrichten dekodiert. Ich deaktiviere jetzt noch die zweite Stelle mit diesen Ausgaben und flashe damit beide SIGNALduinos.

elektron-bbs, 2019-01-18:
Die letzten 24 h gab es keine Auffälligkeiten, beide SIGNALduino lieferten normale Anzahlen an Nachrichten.
Seit etwa einer Stunde habe ich die aktuelle Version auf allen 4 SIGNALduinos (1 x NanoCC1101, 1 x NanoCC1100, 1 x Nano, 1 x RadinoCC1101). Bis jetzt sieht es gut aus.

sidey79, 2019-01-18:
Es würde mich insgesamt sehr freuen, wenn wir den Fehler damit identifiziert hätten.
Zum einen, weil er dann nur durch das erweiterte Debugging zustande kam und zum anderen, weil Du ihn recht schnell identifizieren konntest.

Danke für die tatkräftige Unterstützung

sidey79, 2019-01-20:
@elektron-bbs 

Was meinst Du. Können wir den Fall schließen oder hast Du noch Zweifel?

elektron-bbs, 2019-01-20:
Meine SIGNALduinos machen derzeit seit etwa 2 d Uptime keine Probleme mehr.

closed 2019-01-20 by elektron-bbs