RFD-FHEM/SIGNALDuino #111: Besserer Name für nighty Release

sidey79, 2019-02-11:


sidey79, 2019-02-11:
@elektron-bbs @Ralf9 

So hier, wäre der Platz um den Namen des "nightly" Tags zu überdenken.

HomeAutoUser, 2019-02-11:
Wie wäre es mit 
`prerelease`

elektron-bbs, 2019-02-11:
built_2019-02-11

sidey79, 2019-02-11:
Ein Datum fände ich nicht so gut, weil das Tag dann jeden Tag auch einen neuen Namen hat.
Ein gleichbleibender Name wäre da besser, da weniger Aufwand

sidey79, 2019-02-11:
bezüglich prerelease ist das nur ein anderer Begriff für Releasecandidate.
Das passt dann vielleicht nicht so ganz.

Passend wäre vielleicht:
PerpetualBeta

Ralf9, 2019-02-11:
Was spricht dagegen dem Namen das Datum dranzuhängen der Aufbau des Namens wäre dann `Tag_Datum` 
In der `SIGNALduino_githubParseHttpResponse ` wird dann 
`$releaselist.=$item->{name}."," ; `

Ralf9, 2019-02-11:
Bei set flash kann das Datum dann wieder entfernt werden:
```
		my ($tags, undef) = split("_", $args[0]);
		my $ghurl = "https://api.github.com/repos/RFD-FHEM/<REPONAME>/releases/tags/$tags";
```

sidey79, 2019-02-11:
Also dann nightly als Name lassen und vom asset das Feld 'created_at' anhängen?

Würde dann zu

`nightly_2019-02-03T21:37:58Z 
`
führen.

Ralf9, 2019-02-11:
Ja, wäre auch ok

sidey79, 2019-02-11:
Wird dann halt recht lang und bei den anderen Versionen würde das Datum auch dran stehen

Ralf9, 2019-02-12:
> und vom asset das Feld 'created_at' anhängen

Bei älteren Releases (z.B. 3.3.0) die nachträglich in die Releases Seite hinzugefügt wurden, passt das asset Feld 'created_at' nicht. asset created_at ist das Datum, wenn die hex Dateien hochgeladen wurden.
Das created_at vom Release passt besser.
So wie Du eine neue Nightly Version in die Releases Seite einttägst, wird auch das created_at vom Release aktualisiert.

> Wird dann halt recht lang

Damit wird es nicht so lang
`$releaselist.=$item->{tag_name}."__".substr($item->{created_at},0,10)."," ;`

sidey79, 2019-02-12:
> Bei älteren Releases (z.B. 3.3.0) die nachträglich in die Releases Seite hinzugefügt wurden, passt das asset Feld 'created_at' nicht. asset created_at ist das Datum, wenn die hex Dateien hochgeladen [...]

sidey79, 2019-02-12:
> Bei älteren Releases (z.B. 3.3.0) die nachträglich in die Releases Seite hinzugefügt wurden, passt das asset Feld 'created_at' nicht. asset created_at ist das Datum, wenn die hex Dateien hochgeladen [...]

Sieht seltsam aus, dass es vor dem Erstellen gepublisht wurde.  Aber es ist wohl so.
```
  "created_at": "2019-02-03T21:15:30Z",
  "published_at": "2019-01-04T22:57:23Z",
```

Wieso hängen wir es nicht ausschließlich bei nightly oder wie auch immer wir es nennen an? Bei den normalen Releases wird das ja eher nicht benötigt denke ich.

Ralf9, 2019-02-12:
Es schadet aber auch nicht, wenn es auch bei den normalen Releases angehängt wird.
Es ist z.B. interessant zu sehen wie alt z.B. das 3.3.0 ist.
Es ist damit auch egal ob es nightly oder anders wie z.B. built heisst

closed 2022-07-05 by elektron-bbs