RFD-FHEM/SIGNALDuino #112: radinocc1101 - V 3.3.1-RC-nightly - closed - only power on,off to reset

HomeAutoUser, 2019-02-12:
Hallo, ich flashte den radino auf die aktuelle nightly Version und seither spinnt er herum.
Dauern wird er geschlossen und ich bin gezwungen diesen vom USB zu trennen um ihn wieder zu reaktivieren.

version | V 3.3.1-RC-nightly SIGNALduino cc1101 (433 Mhz ) - compiled at Feb  3 2019 21:35:28

```
2019.02.13 00:25:08 3: radino_433Mhz device opened
2019.02.13 00:25:10 3: radino_433Mhz/init: disable receiver (XQ)
2019.02.13 00:25:10 3: radino_433Mhz/init: get version, retry = 0
2019.02.13 00:25:20 3: radino_433Mhz/init: get version, retry = 1
2019.02.13 00:25:30 3: radino_433Mhz/init: get version, retry = 2
2019.02.13 00:25:40 3: radino_433Mhz/init: get version, retry = 3
2019.02.13 00:25:40 2: radino_433Mhz/init retry count reached. Closed
2019.02.13 00:25:40 2: radino_433Mhz closed
```

HomeAutoUser, 2019-02-13:
@sidey79, kann es sein, das im derzeit hinterlegten HEX file ein Fehler ist oder nicht der richtige Prozessor ausgewählt oder ...

sidey79, 2019-02-13:
Ich werde es auf einen armega32u4 flashen, dann wissen wir mehr.

sidey79, 2019-02-13:
Habe die Radino Firmware auf einen atmega32u4 gebracht, der denkt ein radino zu sein.

Also die Firmware funktioniert. Allerdings ist in der Firmware eine Baudrate von 57600 compiliert.
Waren dass die alten nicht auch?

HomeAutoUser, 2019-02-13:
Ich denke sie waren es.

Da ich heute das Phänomen wieder hatte, habe ich eine Vermutung. Vermutlich kann ich den radino mit einem anderen Empfänger "abschießen" wenn ich etwas längeres sende. Dem muss ich aber erst nochmal morgen auf die Spur gehen.

sidey79, 2019-02-14:
Hast Du die Baudrate in deiner Definition angepasst?

elektron-bbs, 2019-02-14:
Die Baudrate ist doch schon immer 57600.

HomeAutoUser, 2020-01-13:
Problem behoben

closed 2020-01-13 by HomeAutoUser