RFD-FHEM/SIGNALDuino #113: Zuverlässiger Support für TFA 30.3200

cxxcoder, 2019-03-19:
Hallo,

ich besitze mehrere Sensoren des Typs TFA 30.3200, die auch mal bei Conrad unter der Typbezeichung S014 verkauft wurden, und verwende einen nanoCC1101. Standardmäßig wurde jedoch nur einer der Sensoren als Protokoll 33 erkannt. Ich habe aber herausgefunden, indem ich die Rohdaten aufgezeichnet habe, dass mit geänderten Timings alle empfangenen Nachrichten zuverlässig erkannt werden:

`
-                       one                                             => [1,-8],
-                       zero                                    => [1,-4],
-                       sync                                    => [1,-16],
-                       clockabs                        => '500',
+                       one                                             => [1,-5.6],
+                       zero                                    => [1,-2.8],
+                       sync                                    => [1,-11],
+                       clockabs                        => '736',
`

Diese krummen Werte habe ich als Mittelwerte aus den Rohdaten bestimmt.
Zunächst einmal stellt sich die Frage, ob diese Sensoren ebenfalls von Protokoll 33 erfasst werden sollten oder ob es eigentlich ein anderes Protokoll ist, das nur ähnlich ist. Ich könnte auch eine CRC Methode für diese Sensoren beitragen, die beim Protokoll 33 noch fehlt.

Allerdings habe ich dann leider festgestellt, dass bereits in den Rohdaten nicht alle Nachrichten der Sensoren zuverlässig empfangen werden, es also auch SIGNALDuino-seitig ein Problem mit den Sensoren gibt. Mal wird ein Sensor zuverlässig für ein paar Stunden empfangen, dann mal wieder für ein paar Stunden nicht. Ich tippe daher funkseitig auf ein Sync-Problem, da eigentlich die RSSI-Werte, wenn empfangen wird, OK sind. Vielleicht gibt es da auch einen Zusammenhang mit den krummen Timing-Werten. Funktechnisch bin ich da leider kein Experte.

Getestet habe ich mit 3.3.1-RC4 und 3.3.1-RC10. Verhalten war beides Mal das gleiche. Ich habe es aber auch hinbekommen, SIGNALDuino selbst zu bauen. Wenn jemand etwas damit anfangen kann, könnte ich auch mal eine Debug-Version bauen und die Ausgabe mitschneiden.

Ich wäre aber auch bereit, mich selbst einzuarbeiten und etwas zu Debuggen. Allerdings konnte ich den Code in der signalDecoder.cpp dazu bisher nicht ausreichend nachvollziehen. Wenn es irgendwo weiterführende Dokumentation gibt, die bei der Einarbeitung helfen würde, wäre ich für einen Hinweis dankbar.

HomeAutoUser, 2019-03-19:
Guten Tag @cxxcoder,
danke für deine Meldung.

- Kannst du bitte mal beschreiben wie du die Zeiten aufgenommen bzw. aufgezeichnet hast.
- Es wäre sehr hilfreich wenn du mal ein paar RAWMSG sammeln würdest von dem Sensor mit der aktuellen Firmware bitte.

Die Protokolldefinition welche du in der Datei vorgefunden hast, ist eine "Rundung" welche damals von einigen Nachrichten vorgenommen wurde. Da ich von der Protokoll die Sensoren renkforce E0001PA, Conrad S522 und TX-EZ6 untersucht habe, so wäre es SEHR hilfreich wenn du deine Timings bestmöglich vielleicht mit einem LogicAnalyser messen könntest.

Die einzelenen Protokolldefinitionen müssen nicht immer exakt sein, da auch eine Tolleranz einspielt. Um das Ganze aber bewerten zu können, so benötigen wir mehr Input von dir.

- CRC Methode, dafür sind wir immer offen wenn du die Berechnung uns mitteilen könntest.

Mit mehr Informationen, wird bestimmt auch @elektron-bbs mal über den Sachverhalt mit blicken, der das Protokoll letztens in der mache hatte. (TX-EZ6 (Weatherstation TZS First Austria)

MfG

elektron-bbs, 2019-03-19:
Ein clockabs von 736 wäre schon eine ziemlich heftige Abweichung. Bei pilight https://manual.pilight.org/protocols/433.92/weather/dostmann.html sind sie auf 500 bzw. 600 gekommen. Hier https://github.com/RFD-FHEM/RFFHEM/issues/474 hatten wir einen Sensor mit ca. 400 und hier wurde der S014 https://github.com/RFD-FHEM/RFFHEM/issues/53 mit 500 angelegt.
Evtl. müssten wir zwei neue Definitionen hinzufügen. Allerdings sind auch die Einstellungen des CC1101 mitverantwortlich für die Timings. Bitte poste mal deine Einstellungen.

Bezüglich des Prüfverfahrens kannst du dich ja schonmal hieran versuchen :-)
[SD_WS_33.xlsx](https://github.com/RFD-FHEM/SIGNALDuino/files/2984552/SD_WS_33.xlsx)

cxxcoder, 2019-03-19:
Hallo,

in FHEM habe ich für den SIGNALDuino folgende Einstellungen bzw. Readings:

```
ccconf		freq:433.920MHz bWidth:325KHz rAmpl:42dB sens:4dB (DataRate:5603.79Baud)
ccpatable	C3E = 00 84 00 00 00 00 00 00 => 5_dBm
```

Gibt's da noch irgendetwas anderes, was eingestellt werden muss? Wäre natürlich die einfachste Erklärung, dass der CC1101 falsch konfiguriert ist, insbesondere für die stundenweisen Aussetzer beim Empfang.

Ich habe mal 25 Minuten die Raw Messages mitgeschnitten, siehe anbei. Da ich 5 Sensoren diese Typs im Einsatz habe und sie jede Minute ein Telegramm senden, kommt da schon etwas zusammen. Die passende CRC Methode lautet:

```perl
crcok => sub {
	my $crc = 0;
	for (my $i=0; $i < 34; $i++) {
		if (substr($bitData, $i, 1) == ($crc & 1)) {
		   $crc = ($crc>>1);
		} else {
		   $crc = ($crc>>1) ^ 12;
		}
	}
	$crc ^= SD_WS_bin2dec(reverse(substr($bitData, 34, 4)));
	return $crc == SD_WS_bin2dec(reverse(substr($bitData, 38, 4))) ? 1 : 0;
},
```

Über einen LogicAnalyser verfüge ich leider nicht. Das einzige, was ich anbieten kann, ist, eine spezielle Debug-Version auf den nanoCC1101 zu flashen und deren Output zu dumpen.
Übrigens lassen sich die Sensoren mit pilight und einem dieser China OOK Empfänger am GPIO absolut zuverlässig empfangen. Ich würde jedoch gerne pilight in FHEM loswerden und gerne vollständig durch SIGNALDuino ersetzen. Falls dies weiterhilft, könnte ich auch noch pilight-debug dumps der Sensoren anbieten oder in irgendeiner anderen Form etwas über den China OOK Empfänger dumpen.

MfG

[sd_raw.zip](https://github.com/RFD-FHEM/SIGNALDuino/files/2984626/sd_raw.zip)

elektron-bbs, 2019-03-19:
Ich bevorzuge für ccconf sens:8. Das dürfte allerdings nicht allzu viel ausmachen.

Die Prüfung musste ich ein klein wenig ändern, dann passt es auch bei meinen Sensoren (3 x S522) und beim Dispatchen der Nachrichten deiner Sensoren:

											$crc ^= SD_WS_bin2dec(reverse(substr($bitData, 34, 4)));
											if ($crc - SD_WS_bin2dec(reverse(substr($bitData, 38, 4))) == 0) {
												return 1;
											} else {
												Log3 $name, 3, "$name: SD_WS_33 Parse msg $msg - ERROR check $crc != " . SD_WS_bin2dec(reverse(substr($bitData, 38, 4)));
												return 0;
											}
											# return $crc == SD_WS_bin2dec(reverse(substr($bitData, 38, 4))) ? 1 : 0;

cxxcoder, 2019-03-19:
Einen Temperatur- oder Feuchtetrend enthalten die Telegramme meines Wissens nach nicht.
Ehrlich gesagt würde ich erst einmal gerne damit weiterkommen, dass der Empfang zuverlässig wird und ich nicht mehr stundenweise Aussetzer habe, denn damit ist für mich SIGNALDuino für diese Sensoren unbrauchbar.
Hat hier keiner sonst diesen Sensor und kann zumindest checken, ob er vergleichbare raw messages bekommt? Sie werden ja auf der [FHEM Seite](https://wiki.fhem.de/wiki/SIGNALduino) explizit genannt. Vielleicht stimmt auch etwas mit meinem nanoCC1101 nicht.

Ralf9, 2019-03-19:
Wenn der China OOK Empfänger ein rxb6 ist, dann kannst du mit einem arduino nano damit auch einen Signalduino aufbauen.
Evtl bringt es auch was beim cc001 die bwith zu erhöhen

cxxcoder, 2019-03-20:
Auch mit maximaler bandwidth ist der Empfang leider nicht zuverlässig. Den CC1101 würde ich schon gerne verwenden, da ich auch einen TFA Sensor habe, der nicht OOK verwendet und daher vom RXB6 nicht empfangen werden kann.
Also falls einer den TFA 30.3200 mit CC11010 verwendet und zuverlässigen Empfang hat, wäre ich dankbar für einen Hinweis, welche Hard- und Firmware genau verwendet wird. Im Internet findet man ja mehrere Varianten des CC1101 für 433MHz.

elektron-bbs, 2019-03-20:
Mhmm, das verstehe ich jetzt nicht. Wie hast du den CC1101 konfiguriert, das er zum einen einen Sensor, der nicht OOK verwendet, empfängt und zum anderen den TFA30.3200 mit OOK empfangen soll?

Wie äußern sich eigentlich die Aussetzer? Hast du da plötzlich gar keinen Empfang mehr, oder setzt nur ein oder alle Sensoren vom gleichen Typ aus?
Funktioniert der Empfang dann genau so plötzlich ohne einen Eingriff von allein wieder?
Irgendwelche Störquellen im Haus? Es wrde auch schon von Aktoren berichtet, die plötzlich zum Dauersender wurden. Oder auch Sensoren, die wegen schwacher Batterie dann Dauerstrich senden und somit sämtlichen Funkverkehr blockieren.
Stromversorgung des SIGNALduino kontrolliert?

cxxcoder, 2019-03-20:
Ich dachte, der CC1101 könnte mehrere Modulationsarten parallel empfangen. Zumindest habe ich noch einen TFA 30.5045.54 und, da dessen Signale von pilight mit dem OOK Empfänger nicht empfangen wurden, bin ich davon ausgegangen, dass er kein OOK verwendet. Kann aber natürlich auch einen anderen Grund haben.
Zumindest kann SIGNALDuino diesen Sensor sowie den TFA 30.3200 empfangen, aber den TFA 30.3200 bei mir nicht zuverlässig. Meine Hoffnung war eben, dass ich beide Sensoren mit SIGNALDuino zuverlässig empfangen kann.
SIGNALDuino hängt direkt am Raspberry. Ich könnte natürlich jetzt noch verschiedene Dinge durchprobieren: Netzteil wechseln, Kabel wechseln, Standort wechseln etc.
Was mich jedoch stutzig gemacht hat, ist, das die 30.3200-er mal für ein paar Stunden empfangen werden und mal für ein paar Stunden nicht und mal der eine empfangen wird und der andere nicht und dann wieder umgekehrt. Der Empfang kommt auch wieder, ohne das ich etwas unternehmen muss. Ehrlich gesagt wüsste ich nicht, wie sich dieses Verhalten durch Störquellen oder Probleme mit der Stromversorgung erklären lassen sollte, denn dann sollten ja immer alle Sensoren gleich betroffen sein und nicht mal der eine und mal der andere.
Meine Idee war daher, dass vielleicht ein Sync-Problem vorliegt und es zu einem Art Schwebungseffekt kommt. Vielleicht ist es die einfachste Lösung, die 30.3200 durch 30.5045.54 zu ersetzen.

elektron-bbs, 2019-03-20:
Eine Möglichkeit wäre noch, das der SIGNALduino die Empfangspakete manchmal nicht als MS auswertet und sie stattdessen als MU-Nachricht ausgibt. Wir haben einige Protokolle, wo wir eine Definition mit MS und eine für MU anlegen mussten, weil sich der SIGNALduino manchmal nicht entscheiden kann. Du könnest mal probeweise eine zweite Definition einbauen, gleich der für Protokoll 33 und ersetzt dort "sync" durch "start", wie z.B. bei folgenden Protokollen:
13 und 13.1
72 und 72.1
91 und 91.1

cxxcoder, 2019-03-20:
Ich habe zum Testen die 00_SIGNALduino.pm angepasst und direkt die Raw-Pakete mitgeloggt. Wenn ein TFA 30.3200 nicht empfangen wird, kommen zu der Zeit, wenn er sendet, keinerlei Raw-Pakete an, auch nicht MU.
Ich habe auch schon mal in den SIGNALDuino Code reingeschaut, aber ohne etwas Dokumentation und Beschreibung, was da passiert, ist das wenig verständlich. Interessant wäre natürlich, in irgendeiner Form die Rohdaten des CC1101 aufzuzeichnen und zu checken, ob darin die Telegramme erkennbar sind. Falls ja, würde dies auf einen Bug im Arduino-Code hindeuten. Vielleicht probiere ich es mal, wenn ich Zeit und Muse habe.

elektron-bbs, 2019-04-19:
Die Protokolldefinition und das Prüfverfahren wurden jetzt in die Entwicklerversion eingearbeitet.

elektron-bbs, 2019-06-20:
Hallo @cxxcoder wie ist jetzt eigentlich hier der Stand? Ist das Problem gelöst?

cxxcoder, 2019-06-24:
Hallo @elektron-bbs, danke der Nachfrage. Das Problem ist noch nicht gelöst, aber es kann sein, dass es an meinem SIGNALduino liegt. Letztlich wird kein Weg daran vorbeiführen, sich die Rohdaten, die vom CC1101 Chip kommen, anzuschauen. Leider hatte ich bisher keine Zeit, dies zu tun.

Ralf9, 2019-06-24:
Hast Du es auch schon mal mit deaktiviertem MS- und MC-Decoder versucht?
`config  MS=0;MU=1;MC=0;...`

elektron-bbs, 2022-07-05:
Da es hier seit drei Jahren keine Reaktion mehr gibt, würde ich vorschlagen, das Issue zu schließen.

closed 2022-07-06 by cxxcoder