RFD-FHEM/SIGNALDuino #114: Fehler bei MU-Dekodierung

elektron-bbs, 2019-05-06:
Ausgehend von diesem Issue [ME Funkklingel](https://github.com/RFD-FHEM/RFFHEM/issues/367#issuecomment-486769322) haben wir einen Fehler in der Firmware des SIGNALduino entdeckt. Bei diesem Protokoll (79) wird nur jede zweite Nachricht dekodiert. Um das nachvollziehen zu können, habe ich mir ein kleines Sendeprogramm geschrieben, das mit MSG 0x610 beginnend sendet und dann fortlaufend bis 0x61F die Sendungen weiterführt.
[SIGNALduino-Send_Heidemann_610-61F.zip](https://github.com/RFD-FHEM/SIGNALDuino/files/3149382/SIGNALduino-Send_Heidemann_610-61F.zip)

Dekodiert werden nur MSG, die in diesem Fall ungerade sind, also 611, 613, 615 u.s.w.
[image]

Bei abgeschaltetem MS (MS=0;MU=1;MC=1;Mred=1) werden alle MSG dekodiert.

Getestet wurde mit verschiedenen SIGNALduinos:
Radino mit CC1101, Nano mit CC1101 und ohne, SIGNALESP mit CC1101
jeweils mit aktueller Firm- und Software:
version: V 3.3.1-RC-nightly_2019-02-05 SIGNALduino cc1101 (chip CC1101) 
versionProtocols: 1.03 
versionmodul: v3.4.0_dev_16.03 

Ein ausführliches Log mit verbose 5 hänge ich an.
[Empfangstest_Heidemann_610-61F_fhem-2019-05-06.log](https://github.com/RFD-FHEM/SIGNALDuino/files/3149413/Empfangstest_Heidemann_610-61F_fhem-2019-05-06.log)

HomeAutoUser, 2019-05-11:
@sidey79 auf das Problem möchte ich dich auch nochmal hinweisen weil dies auch Priorität haben sollte ;-)
Wenn du in einer Art Hilfe benötigst bzw. uns Punkte zum ansetzen nennen könntest, so kann man dir helfen. Leider wissen wir nicht wie die decodierung im Grunde komplett funktioniert.

HomeAutoUser, 2019-05-19:
@sidey79 ,
ich muss noch einmal hier angreifen um das Problem zu erwähnen ;) oder bist du diesbezüglich ratlos?

Diesen gravierenden Unterschied der Erkennung / Bug würde ich sehr gern gleich stellen wollen mit Ralfs FW da wir nicht sicherstellen können, das es bei anderen Sensoren ebenso ist.

sidey79, 2019-05-19:
Ich hatte das Problem gedanklich mit einem anderen in Verbindung gebracht und es daher nicht weiter verfolgt.

Sorry. Ist mir durch die Lappen gegangen :(
Ich werde es analysieren

sidey79, 2019-05-23:
Kuzer Zwischenstand:

Die Nachrichten werden als MS Nachrichten erkannt. Clock und Sync Sequenz passen.
Da die Nachricht aber nur 25 Pulse beinhaltet, wird die Nachricht verworfen und es kommt zu keiner Ausgabe.

HomeAutoUser, 2019-06-03:
Gibt es hier schon neue Erkenntnisse?

sidey79, 2019-06-04:
Ja, es handelt sich um eine MS Signal.

Allerdings wäre hier noch zu klären, ob das Prüfverfahren hier in Ordnung ist.

elektron-bbs, 2019-06-04:
Die SIGNALduinos werfen aber nur MU aus, wenn das letzte Bit eine 1 ist. Wenn das letzte Bit eine 0 ist, kommt zu 99% gar keine Message vom SIGNALduino.

sidey79, 2019-06-04:
Das konnte ich bislang nicht nachvollziehen, dass es etwas mit dem letzten Bit zu tun hat. Wie hast Du das festgemacht?

Die Ausgabe als MS habe ich schon herbeigeführt, allerdings würde ich gerne mal verstehen, ob diese Klingel tatsächlich mehrere Übertragungen mit unterschiedlichen Signalen vor nimmt.

elektron-bbs, 2019-06-04:
Es sind keine unterschiedlichen Signale. Die Nachrichten unterscheiden sich nur im letzten Bit. Ich dachte eigentlich, ich hätte das im ursprünglichen Post gut genug dokumentiert.

sidey79, 2019-06-04:
Ja, das Vorgehen in deinem Testprogramm ist doch, dass Du 5x Nachricht A, dann 5x Nachricht B, usw. sendest.

Die Klingel sendet aber immer nur 5x Nachricht A und macht dann Schluss?

HomeAutoUser, 2019-06-05:
>Das konnte ich bislang nicht nachvollziehen, dass es etwas mit dem letzten Bit zu tun hat. Wie hast Du das festgemacht?

Diese Erkenntnis sieht man, wenn das Testprogramm das letzte Bit ändert und ein neuer Code gesendet wird. Das Testprogramm simuliert sozusagen die Klingel mit verschiedenen Codes.

Hier https://github.com/RFD-FHEM/SIGNALDuino/issues/114#issue-440838157 siehst du, das die Decodierung von Dir nur ungereade Codes erkennt obwohl auch gerade Codes gesendet werden. Die Codes wie 612 614 616 unsw. werden von DEINER Firmware NICHT erkannt, **dies IST DER BUG !**

elektron-bbs, 2019-06-05:
Ich zitiere mal noch eine Zusammenfassung des Users @Holger-Arnold in [ME Funkklingel](https://github.com/RFD-FHEM/RFFHEM/issues/367#issuecomment-487382776):

"Nach dem Senden der 8 möglichen Melodien (mit dem Sender 70372) bekomme ich reproduzierbar 3 davon als SD_BELL durch den Signalduino angelernt.
Es sind immer wieder dieselben, nämlich die mit den Endungen
SD_BELL_xx1, SD_BELL_xx9, SD_BELL_xxD."

"Die 5 nicht automatisch angelernten Melodien habe ich durch Ausprobieren herausgefunden (alles zwischen xx0..xxF manuell hinzugefügt und hinreichend oft gesendet)"

Dieses Verhalten habe ich mit dem Testsender reproduziert, halt nur mit allen 16 Möglichkeiten.

Holger-Arnold, 2019-06-05:
Korrekt. Weitere Hinweise aus dem anderen Thread:
- 25. April 2019: 'Gut-Fall' der 3 funktionierenden Melodien geloggt, vielleicht hilft das bei der Analyse, wie sich der Sender verhält.
- 27. April: mit <10cm funktioniert der Empfang für alle 8 Melodien
- 27. April: Ohne MS- und MC-Decoder funktioniert der Empfang für alle 8 Melodien

sidey79, 2019-06-06:
Ich denke wir müssen mit einer angepassten Firmware und angepasster Protokoll Definition  noch mal testen.

Den anderen Fall habe ich verstanden.
Wenn 20x 0x610 gesendet wird, dann wird es als MU ausgegeben und wenn 20x 0x611 gesendet wird, dann geht es nicht .

HomeAutoUser, 2019-06-06:
> Den anderen Fall habe ich verstanden.
Wenn 20x 0x610 gesendet wird, dann wird es als MU ausgegeben und wenn 20x 0x611 gesendet wird, dann geht es nicht .

Genau richtig.

Es ist auf jedenfall ein Unterschied in der Verarbeitung bei dir und @Ralf9. Vielleicht kann er Dir einen Tip geben wo vielleicht.

Solche Differenzen sollten wir beheben und wenn „erledigt“ dann ein SVN Update vornehmen weil sonst ereilen uns diese „Schleichfehler“ von Usern noch länger.

Ralf9, 2019-06-06:
die notwendige Anpassung ist eigentlich recht einfach, es muss nur geprüft werden ob die empfangene MS-Nachricht kürzer als die minMessageLen ist und dann als MU-Nachricht ausgegeben werden.

HomeAutoUser, 2019-06-07:
** vorsichtig anfrag**

@sidey79, wie schätzt du diesen Bug ein um dies zu beheben? Ich möchte Dich nicht unter Druck setzen aber ich habe noch ein paar andere Differenzen welche ich "offenlegen" würde aber dazu ist es vermutlich ratsam Bug für Bug zu beheben ehe ich "dich zupflastere".

Ich mache mir soeben die Mühe und vergleiche sämtliche Auswertungen von deiner FW mit @Ralf9 seiner und sehe mich und @elektron-bbs jetzt schon bestätigt, ES MUSS unbedingt die Gleichheit hergestellt werden. Es kann nicht sein, das bei dir was anderes raus kommt als bei @Ralf9.

Für jede Sache würde ich einen ISSUES anlegen.

sidey79, 2019-06-07:
Ich bin dran. Die Mindestlänge anpassen ist nicht das Problem. Das habe ich schon gemacht .

Den Fehler mit der fehlenden MU Ausgabe habe ich bislang nicht reproduziert bekommen.
Wenn der MS Decoder deaktiviert ist, dann wird jede Nachricht als MU ausgegeben. Egal ob 5x 0x610 oder 5x 0x611 gesendet wird.

Aktuell bin ich dabei die Auswirkungen dieser Anpassung auf andere Tests zu bewerten. Das sind einige, die jetzt nicht mehr ihr OK melden.

HomeAutoUser, 2019-06-07:
Wenn du den Fehler gefixst hast, so würde ich die Aufzeichnungen und Fehler von mir erneut testen und wenn diese noch "aktiv sind" so ginge es da gleich weiter.

Wenn wir diese durchgearbeitet haben, so würde ich gern ein SVN update anstreben ;-)

sidey79, 2019-06-07:
SVN Update klingt immer gut, was genau soll da aktualisiert werden?

HomeAutoUser, 2019-06-07:
>SVN Update klingt immer gut, was genau soll da aktualisiert werden?

Die Fehler in der Decodierung welcher wir fixen + im selben Atemzug die Module wo sich was änderte.

HomeAutoUser, 2019-06-10:
>die notwendige Anpassung ist eigentlich recht einfach, es muss nur geprüft werden ob die empfangene MS-Nachricht kürzer als die minMessageLen ist und dann als MU-Nachricht ausgegeben werden.

Ich muss nun mal zum Verständnis bitte nachfragen.
@Ralf9, lese ich das richtig heraus, das du das Signal nur als MU verarbeitest und @sidey79 sich zusätzlich dem MS Signal stellt mit der neuen Definition?

sidey79, 2019-06-10:
@HomeAutoUser 

Mein 1. Eindruck war, dass es ein MS Signal ist.
So ganz stimmt das aber nicht, Es wird nur passend im Puffer abgelegt, wenn das letzte Bit den passenden Zustand liefert.

Ich nehme an, dass man das Signal auch als MS erkennen kann. Mittels angepasstem Sync (nur ein Puls) und ReconstructBit  würde man mit hoher Wahrscheinlichkeit das korrekte Ergebnis erhalten.
Würde das Signal aus 40 Pulsen bestehen, hätte man es so auch hinnehmen müssen, dass es ein MS Signal ist. Zumindest passen die Merkmale. Es ist nur zu kurz.

Ich bin jetzt aber noch mal auf die MU Variante umgestiegen, da das Erkennen davon abhängt, welches Bit das letzte einer Übertragung ist.

`MU;P0=-92;P1=364;P2=-676;P4=-327;P5=680;P6=-5240;D=01214545212121214521212145612145452121212145212121456121454521212121452121214561214545212121214521212145612145452121212145212121456121454521212121452121214561214545212121214521212145612145452121212145212121456121454521212121452121214561214545212121214521;CP=1;R=25;O;`

HomeAutoUser, 2019-06-10:
>Ich bin jetzt aber noch mal auf die MU Variante umgestiegen, da das Erkennen davon abhängt, welches Bit das letzte einer Übertragung ist.

hieße das, das die Definition MS 79.1 überfällig wäre?

sidey79, 2019-06-10:
Ja, die 79.1 wird dann nicht benötigt.

PR #115 (2019-10-08): Lowered minMessageLen to allow MS decoder detecting shorter ms signals
Lowered minMessageLen to fix #114

Added testcase muHeidemann

Added testcase msHeidemann

Todo:
Check failing tests
commit a2de913d (2019-10-08): Lowered minMessageLen to allow MS decoder detecting shorter ms signals (#115)

* SignalDetectorClass
- Lowered minMessageLen to fix #114
- If MS message is to short skip MS processing and go allow MC/MU processMessage
- Added capability to have more than one sync sequence direct at beginning of the signal. This fix will output all the syncs in front of the message    fixes #116

* testSignalDecoder/tests.cpp

- Added testcase muHeidemann
- Added testcase msHeidemann
- Added test to verify with data from #116
- Fixed not null terminated string passed to atoi call

* Added Submodules correctly

 [... cut]
closed 2019-12-03 by sidey79