RFD-FHEM/SIGNALDuino #116: ID 33 (MS) decode as MU right or other new?

HomeAutoUser, 2019-06-15:
Hallo, ich habe beim Senior im "Dschungel" einiges endeckt und es wäre gut, wenn Ihr mal drüber schaut. Es gibt immer periodisch aller paar Minuten solche MU Nachrichten welche ich mal ans SD_WS weitergeleitet habe weil die Zeiten der ID 33 entsprechen. Vermutlich passt aber die CRC oder ein Test nicht OBWOHL die Logausgabe FEHLER bei Log3 erfolgen sollte.

```
		"33.3"	=>	## Thermo-/Hygrosensor S014, renkforce E0001PA, Conrad S522, TX-EZ6 (Weatherstation TZS First Austria)
				# ID 33 decode as MU !!!
				# MU;P0=-2908;P1=488;P2=-8144;P3=-2085;P4=-4160;D=01212121212121212121212121212131314131314141314141313131414141313131414141314141313131414131314141313131313131313121213131413131414131414131313141414131313141414131414131313141413131414131313131313131312121313141313141413141413131314141413131314141413141;CP=1;R=26;O;
				# MU;P0=-2764;P1=499;P2=-8160;P3=-2088;P4=-4155;D=01212121212121212121212121212131314131314141314141313131414141313131414141314141313131414131314141313131313131313121213131413131414131414131313141414131313141414131414131313141413131414131313131313131312121313141313141413141413131314141413131314141413141;CP=1;R=23;O;
			{
				name					=> 'weather',
				comment				=> 'S014, TFA 30.3200, TCM, Conrad S522, renkforce E0001PA, TX-EZ6 (decode as MU)',
				id              => '33.3',
				knownFreqs      => '433.92',
				one             => [1,-8],		# 500, -4000
				zero            => [1,-4],		# 500, -2000
				start           => [1,-16],		# 500, -8000
				clockabs        => 500,
				format					=> 'twostate',	# not used now
				preamble				=> 'W33#',			# prepend to converted message
				postamble				=> '',					# Append to converted message
				clientmodule		=> 'SD_WS',
				length_min    => '42',
				length_max    => '44',
			},
```

```
2019.06.14 20:31:27 4: sduino_433/msg READredu: MU;P0=-1188;P1=500;P2=-8141;P3=-2055;P4=-4148;D=01212121212121212121212121212131314131314141314141314131413141314131414141314141313141414131314141313131314131413121213131413131414131414131413141314131413141414131414131314141413131414131313131413141312121313141313141413141413141314131413141314141413141;CP=1;R=20;O;
2019.06.14 20:31:27 4: sduino_433: Fingerprint for MU Protocol id 33.3 -> unbekannt2 matches, trying to demodulate
2019.06.14 20:31:27 4: sduino_433: Decoded matched MU Protocol id 33.3 dmsg W33#26D55D9CC28 length 44 dispatch(1/4) RSSI = -64
2019.06.14 20:56:02 4: sduino_433/msg READredu: MU;P0=-4175;P1=100;P2=-828;P4=-21102;P5=493;P6=-8143;P7=-2086;D=12141456565656565656565656565656565757505757505057505057575750575057575750505057505057505757575757505057575757575057575656575750575750505750505757575057505757575050505750505750575757575750505757575757505757565657575057575050575050575757505750575757505050;CP=5;R=26;O;
2019.06.14 20:56:02 4: sduino_433: Fingerprint for MU Protocol id 33.3 -> weather matches, trying to demodulate
2019.06.14 20:56:02 4: sduino_433: Decoded matched MU Protocol id 33.3 dmsg W33#26C51DA0C10 length 44 dispatch(1/4) RSSI = -61
2019.06.14 20:57:49 4: sduino_433/msg READredu: MU;P0=487;P1=-4156;P2=-2103;P3=-8150;D=01010201020202020201010202020202010202030302020102020101020101020202010201020202010101020101020102020202020101020202020201020203030202010202010102010102020201020102020201010102010102010202020202010102020202020102020303020201020201010201010202020102010202;CP=0;R=27;O;
2019.06.14 20:57:49 4: sduino_433: Fingerprint for MU Protocol id 33.3 -> weather matches, trying to demodulate
2019.06.14 20:57:49 4: sduino_433: Decoded matched MU Protocol id 33.3 dmsg W33#26C51DA0C10 length 44 dispatch(1/4) RSSI = -60.5
2019.06.14 20:57:49 4: sduino_433 Dispatch: W33#26C51DA0C10, Dropped due to short time or equal msg
2019.06.14 20:57:49 4: sduino_433: Decoded matched MU Protocol id 33.3 dmsg W33#26C51DA0C10 length 44 dispatch(2/4) RSSI = -60.5
2019.06.14 20:57:49 4: sduino_433 Dispatch: W33#26C51DA0C10, Dropped due to short time or equal msg
```

Alternativ hänge ich mal die Aufzeichungen der Bits an, welche anfangs länger als ID 98 mitgeloggt wurden.

[SIGNALduino_unknown_98-2019-06.log](https://github.com/RFD-FHEM/RFFHEM/files/3291841/SIGNALduino_unknown_98-2019-06.log)

version | V 3.3.1-experimental SIGNALduino cc1101 (chip CC1101) - compiled at Jun 10 2019 10:18:02
-- | --
versionProtocols | 1.04
versionmodul | v3.4.0_dev_08.06

elektron-bbs, 2019-06-16:
Ich habe diese Version:
V 3.3.1-experimental SIGNALduino cc1101 (chip CC1101) - compiled at Jun 16 2019 00:11:55
seit etwa 3 Stunden auf 4 SIGNALduinos laufen. Bei mir funktioniert wieder alles.

sidey79, 2019-06-16:
Das freut mich zu hören.
Zumindest weiss ich jetzt wieder, warum ich mich damals entschieden hatte die Daten vor einem Sync zu entfernen, wenn sie kürzer als die minMessageLength waren.
Dadurch wurden die mehrfach vorkommenden Sync mit eleminiert

commit e65e2001 (2019-06-15): signalDecoder/src/signalDecoder.cpp
Added capability to have more than one sync sequence direct at beginning of the signal.
This fix will output all the syncs in front of the message    fixes #116

Added test to verify with data from #116
commit a2de913d (2019-10-08): Lowered minMessageLen to allow MS decoder detecting shorter ms signals (#115)

* SignalDetectorClass
- Lowered minMessageLen to fix #114
- If MS message is to short skip MS processing and go allow MC/MU processMessage
- Added capability to have more than one sync sequence direct at beginning of the signal. This fix will output all the syncs in front of the message    fixes #116

* testSignalDecoder/tests.cpp

- Added testcase muHeidemann
- Added testcase msHeidemann
- Added test to verify with data from #116
- Fixed not null terminated string passed to atoi call

* Added Submodules correctly

 [... cut]
closed 2019-12-03 by sidey79