RFD-FHEM/SIGNALDuino #117: radino - reset and close

HomeAutoUser, 2019-09-14:
Hallo,

ich habe nun folgendes beobachtet. Der radino hat im Laufenden Betrieb mehrfach ein Reset gemacht und irgendwann erfolgt ein closed. Ich bin der Meninung irgendwo hatten wir schonmal den Verdacht das da was sein wird. (Leider finde ich nun den Faden oder die Äußerung nicht mehr)

Ich kann ausschließen, das der Empfänger NICHT nichts mehr empfangen konnte weil in wenigen Metern ein anderer Empfänger auf der selben Frequenz mit empfängt und das nicht wenig ;-)

Ich habe den dunklen Verdacht, das der reset nicht richtig dort funktioniert oder da ein Haken drin ist.

```
2019.09.14 06:24:51 3: radino_433Mhz/KeepAlive not ok, retry = 2 -> get ping
2019.09.14 06:25:51 3: radino_433Mhz/KeepAlive not ok, retry = 3 -> get ping
2019.09.14 06:26:51 3: radino_433Mhz/keepalive not ok, retry count reached. Reset
2019.09.14 06:26:51 3: radino_433Mhz reset
2019.09.14 06:27:22 3: Opening radino_433Mhz device /dev/ttyACM1
2019.09.14 06:27:22 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.14 06:27:22 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.14 06:27:22 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.14 06:27:22 3: radino_433Mhz device opened

2019.09.14 06:27:23 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.14 06:27:24 3: radino_433Mhz/init: get version, retry = 0
2019.09.14 06:27:34 3: radino_433Mhz/init: get version, retry = 1
2019.09.14 06:27:44 3: radino_433Mhz/init: get version, retry = 2
2019.09.14 06:27:54 3: radino_433Mhz/init: get version, retry = 3
2019.09.14 06:27:54 2: radino_433Mhz/init retry count reached. Reset
2019.09.14 06:27:54 3: radino_433Mhz reset
2019.09.14 06:28:24 3: Opening radino_433Mhz device /dev/ttyACM1
2019.09.14 06:28:24 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.14 06:28:24 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.14 06:28:24 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.14 06:28:24 3: radino_433Mhz device opened

2019.09.14 06:28:26 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.14 06:28:26 3: radino_433Mhz/init: get version, retry = 0
2019.09.14 06:28:36 3: radino_433Mhz/init: get version, retry = 1
2019.09.14 06:28:46 3: radino_433Mhz/init: get version, retry = 2
2019.09.14 06:28:56 3: radino_433Mhz/init: get version, retry = 3
2019.09.14 06:28:56 2: radino_433Mhz/init retry count reached. Closed
2019.09.14 06:28:56 2: radino_433Mhz closed
```

Erst ein USB ab und USB dran haben ihn wieder auf open und Empfang gestellt.
Ich habe mir sogar die Mühe gemacht und geschaut ob in dem Zeitraum Sensoren eintragen und ja, im Log erhielt ich Sensorwerte über den 2. Empfänger. Ich rede hier auch von Sensoren welche nur ein paar Meter Luftlinie vom radino OHNE Wand senden.

Ist das Verhalten nun normal? Wir haben auf jedenfall keinen 100% reset den wir ausführen bzw. durch etwas wird der radino beeinflusst um nicht richtig neu zu starten.

MfG

elektron-bbs, 2019-09-14:
Bei dem Radino kann der Reset über FHEM nicht funktionieren.
Bei einem "normalen" Arduino erfolgt der Reset durch einfaches Schließen/Öffnen des COM-Ports.
Beim Radino muss die Schnittstelle kurz auf 1200 bps gesetzt werden, um einen Reset auszulösen (siehe z.B.: https://www.mikrocontroller.net/topic/367913 - "Erzwinge Reset durch öffnen/schließen mit 1200bps auf dem Port COM3").

HomeAutoUser, 2019-09-14:
Dann sollte man vielleicht schauen, das wir das einbauen ;)
Es wäre sehr hilfreich.

Ralf9, 2019-09-14:
Im Laufenden Betrieb sollte normalerweise kein Reset notwendig sein, dies sieht eher nach einem Firmware- oder Hardwareproblem aus.
Ich habe hier einen promini mit RXB6 und einen promini mit cc1101 mit jeweils einer uptime von 89 Tagen.

elektron-bbs, 2019-09-14:
@HomeAutoUser 
Du hattest hier https://github.com/RFD-FHEM/SIGNALDuino/issues/112#issuecomment-463409774 ja schon mal eine Vermutung geäußert.

elektron-bbs, 2019-09-14:
Ich habe mir das Problem mit dem Radino heute mal etwas näher angesehen. Eine Lösung habe ich allerdings noch nicht. Evtl. helfen ja meine Erkentnisse, dargestellt in den Kommentaren ja weiter:

sub
SIGNALduino_ResetDevice($)
{
  my ($hash) = @_;
  my $name = $hash->{NAME};
  my $hardware = AttrVal($name,"hardware",0);

  SIGNALduino_Log3 $hash, 3, "$name reset $hardware"; 
  DevIo_CloseDev($hash);
	if ($hardware eq "radinoCC1101") {
		# The reset is triggered when the Micro's virtual (CDC) serial / COM port is opened at 1200 baud and then closed.
		# When this happens, the processor will reset, breaking the USB connection to the computer (meaning that the virtual serial / COM port will disappear).
		# After the processor resets, the bootloader starts, remaining active for about 8 seconds.
		# The bootloader can also be initiated by pressing the reset button on the Micro.
		# Note that when the board first powers up, it will jump straight to the user sketch, if present, rather than initiating the bootloader.	

		# jetzt auf der Konsole ein:
		# stty -F /dev/ttyACM0 ospeed 1200 ispeed 1200
		# und Radino geht in den Bootloader, jetzt koennte man flashen... oder 8 Sekunden warten und mit FHEM den Port wieder oeffnen
		# folgender Perl-Code funktioniert nicht:
		
		my $def = $hash->{DEF};
		my $dev = $hash->{DeviceName};
		my $baudrate;
		($dev, $baudrate) = split("@", $def);
		$hash->{DEF} = $dev . '@1200';
		$hash->{DeviceName} = $dev . '@1200';
		SIGNALduino_Log3 $hash, 3, "$name reset $hardware " . $hash->{DeviceName}; 
		DevIo_OpenDev($hash, 0, undef, undef);
	  $hash->{DEF} = $def;
	  $hash->{DeviceName} = $def;
		DevIo_CloseDev($hash);
	}
  my $ret = DevIo_OpenDev($hash, 0, "SIGNALduino_DoInit", 'SIGNALduino_Connect');

  return $ret;
}

sidey79, 2019-09-14:
Probier es doch mal damit:

DevIo_OpenDev($hash, 1, undef, undef);

elektron-bbs, 2019-09-15:
Das funktioniert leider auch nicht.
Ich habe es auch schon mit sleep(1) zwischen dem Öffnen und Schließen versucht, weil ich dachte, das vieleicht irgendwelche Timings eine Rolle spielen.

Wenn ich es mit "stty -F ..." ausführe, habe ich diese Ausgaben in syslog:

Sep 15 11:20:59 raspberrypi3 kernel: [173028.118291] usb 1-1.4: USB disconnect, device number 20
Sep 15 11:20:59 raspberrypi3 kernel: [173028.412873] usb 1-1.4: new full-speed USB device number 21 using dwc_otg
Sep 15 11:21:00 raspberrypi3 kernel: [173028.556146] usb 1-1.4: New USB device found, idVendor=1da9, idProduct=002b, bcdDevice= 0.01
Sep 15 11:21:00 raspberrypi3 kernel: [173028.556161] usb 1-1.4: New USB device strings: Mfr=2, Product=1, SerialNumber=0
Sep 15 11:21:00 raspberrypi3 kernel: [173028.556171] usb 1-1.4: Product: radino CC1101 
Sep 15 11:21:00 raspberrypi3 kernel: [173028.556181] usb 1-1.4: Manufacturer: In-Circuit          
Sep 15 11:21:00 raspberrypi3 kernel: [173028.557368] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Sep 15 11:21:00 raspberrypi3 mtp-probe: checking bus 1, device 21: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4"
Sep 15 11:21:00 raspberrypi3 mtp-probe: bus: 1, device: 21 was not an MTP device
Sep 15 11:21:07 raspberrypi3 kernel: [173036.310294] usb 1-1.4: USB disconnect, device number 21
Sep 15 11:21:08 raspberrypi3 kernel: [173036.602916] usb 1-1.4: new full-speed USB device number 22 using dwc_otg
Sep 15 11:21:08 raspberrypi3 kernel: [173036.753888] usb 1-1.4: New USB device found, idVendor=1da9, idProduct=002c, bcdDevice= 1.00
Sep 15 11:21:08 raspberrypi3 kernel: [173036.753903] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 15 11:21:08 raspberrypi3 kernel: [173036.753912] usb 1-1.4: Product: radino CC1101
Sep 15 11:21:08 raspberrypi3 kernel: [173036.753921] usb 1-1.4: Manufacturer: Unknown
Sep 15 11:21:08 raspberrypi3 kernel: [173036.755137] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Sep 15 11:21:08 raspberrypi3 mtp-probe: checking bus 1, device 22: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4"
Sep 15 11:21:08 raspberrypi3 mtp-probe: bus: 1, device: 22 was not an MTP device

Mit "set sduinoACM0 reset" sehe ich nichts im syslog.

sidey79, 2019-09-15:
Es gibt durchaus Hoffnung:

https://github.com/sudar/Arduino-Makefile/pull/71/files

Generiere ein neues Serial Port Objekt und versuche dann das zu schließen. Eventuell musst Du DTR auch noch explizit auf 0 setzen:

```
    my $p = Device::SerialPort->new($dev)
      or die "Unable to open $dev: $!\n";
    $p->baudrate(1200);
    $p->write_settings;
    $p->close;
   

```

elektron-bbs, 2019-09-16:
Ich habe jetzt diverse Varianten durchprobiert - negativ. So wie ich das jetzt in diversen Issues in dem von dir verlinkten Projekt mitbekommen habe, ist es wohl ein Perl-Problem. Wenn ich das richtig sehe, haben sie dort diesen Teil letztendlich in Python umgesetzt:
https://github.com/sudar/Arduino-Makefile/blob/master/bin/ard-reset-arduino

sidey79, 2019-09-17:
Irgendwie dachte ich, dass es in dem refenrenzierten PR darum ging von Python zu Perl zu wechseln.
Angeblich wurde etwas gemerged aber im Master ist immer noch ein Python Script.

Woran das nun liegt ist für mich schleierhaft. Entweder hat es nie funktioniert oder es wurde wieder zurück zu Python gewechselt. :(

Mir fallen folgende Optionen ein:

1. Es geht doch mit Perl (sieht ja aktuell eher nicht danach aus)
2. Wir schreiben ein Pyhton script
3. Direkter stty Aufruf - Klappt dann wohl nicht unter Windows
4. Testen ob avrdude den Reset machen kann

sidey79, 2019-09-17:
BTW, die verlinkte Phyton Variante macht folgnendes:

1. Seriellen Port mit 57600 initialisieren
2. Serielle Schnittstelle schließen
3. Baudrate auf 1200 setzen
4. Serielle Schnittstelle öffnen
5. RTS aktivieren
6. DTR deaktivieren
7. Warten bis die Serielle Schnittstelle wieder aktiv ist

elektron-bbs, 2019-09-18:
Ich habe jetzt zumindest schon mal einen Teilerfolg:

sub SIGNALduino_ResetDevice($)
   DevIo_CloseDev($hash);
   system("stty -F $dev ospeed 1200 ispeed 1200");
   sleep(1);    
Radino macht sofort Neustart.

Entscheidend war das sleep nach dem stty, ohne erfolgt keine Reaktion. Ich probiere das nachher nochmal mit reinem Perl.

elektron-bbs, 2019-09-18:
> Mir fallen folgende Optionen ein:
>     1. Es geht doch mit Perl (sieht ja aktuell eher nicht danach aus)
>     2. Wir schreiben ein Pyhton script
>     3. Direkter stty Aufruf - Klappt dann wohl nicht unter Windows
>     4. Testen ob avrdude den Reset machen kann

1. habe ich jetzt nochmals durchprobiert - gelingt mir nicht.
2. erfordert zusätzliche Installationen, getestet habe ich es nicht.
3. funktioniert (zumindest hier)
4. avrdude müsste die Schnittstelle ja auch erst mal mit 1200 bps öffnen, probieren oder warten, schließen, wieder mit 57600 öffnen und dann flashen. Ich denke, das wird nicht funktionieren.

elektron-bbs, 2019-09-18:
Das Flashen der Firmware des Radino funktioniert mit Variante 3 jetzt hier auch, ohne das ich eine Taste am Radino betätigen muss.
Voraussetzung dafür ist allerdings, das der passende Bootloader darauf ist (reset muss 2x betätigt werden) und der SIGNALduino mit /dev/ACM* definiert ist.

HomeAutoUser, 2019-09-19:
Hallo,

ich würde ausschließen wollen, das das Phänomen bei einem bestimmten Signal auftritt.

```
2019.09.16 22:23:18 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.16 22:23:18 3: radino_433Mhz/init: get version, retry = 0(W37#5E330F7E950)
2019.09.16 22:23:18 3: radino_433Mhz/init: MU;P0=609;P1=-248;P2=237;P3=-611;P4=857;P5=-848;D=01230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101;CP=2;R=1;O;
2019.09.16 22:23:29 3: radino_433Mhz/init: get version, retry = 1(W37#5E330F7E950)
2019.09.16 22:23:29 3: radino_433Mhz/init: MU;P0=609;P1=-248;P2=237;P3=-611;P4=857;P5=-848;D=01230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101;CP=2;R=1;O;
2019.09.16 22:23:39 3: radino_433Mhz/init: get version, retry = 2(W37#5E330F7E950)
2019.09.16 22:23:39 3: radino_433Mhz/init: MU;P0=609;P1=-248;P2=237;P3=-611;P4=857;P5=-848;D=01230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101;CP=2;R=1;O;
2019.09.16 22:23:49 3: radino_433Mhz/init: get version, retry = 3(W37#5E330F7E950)
2019.09.16 22:23:49 3: radino_433Mhz/init: MU;P0=609;P1=-248;P2=237;P3=-611;P4=857;P5=-848;D=01230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101010123010101010101230123230123012301234545454523012301010101232323010123230101232323230101;CP=2;R=1;O;
2019.09.16 22:23:49 2: radino_433Mhz/init retry count reached. Closed
2019.09.16 22:23:49 2: radino_433Mhz closed

2019.09.16 22:43:32 3: radino_433Mhz device opened
2019.09.16 22:43:34 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.16 22:43:34 3: radino_433Mhz/init: get version, retry = 0(500A4D100B200600002500)
2019.09.16 22:43:34 3: radino_433Mhz/init: MC;LL=-2711;LH=3144;SL=-1264;SH=1672;D=2BB9FF4E;C=1465;L=32;R=229;
2019.09.16 22:43:44 3: radino_433Mhz/init: get version, retry = 1(500A4D100B200600002500)
2019.09.16 22:43:44 3: radino_433Mhz/init: MC;LL=-2711;LH=3144;SL=-1264;SH=1672;D=2BB9FF4E;C=1465;L=32;R=229;
2019.09.16 22:43:54 3: radino_433Mhz/init: get version, retry = 2(500A4D100B200600002500)
2019.09.16 22:43:54 3: radino_433Mhz/init: MC;LL=-2711;LH=3144;SL=-1264;SH=1672;D=2BB9FF4E;C=1465;L=32;R=229;
2019.09.16 22:44:04 3: radino_433Mhz/init: get version, retry = 3(500A4D100B200600002500)
2019.09.16 22:44:04 3: radino_433Mhz/init: MC;LL=-2711;LH=3144;SL=-1264;SH=1672;D=2BB9FF4E;C=1465;L=32;R=229;
2019.09.16 22:44:04 2: radino_433Mhz/init retry count reached. Closed
2019.09.16 22:44:04 2: radino_433Mhz closed

2019.09.19 17:24:58 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.19 17:24:58 3: radino_433Mhz/init: get version, retry = 0(W37#5E330F7FF70)
2019.09.19 17:24:58 3: radino_433Mhz/init: MU;P0=-432;P1=849;P2=-861;P3=242;P4=-609;P5=602;P6=-243;D=01212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565;CP=5;R=0;O;
2019.09.19 17:25:08 3: radino_433Mhz/init: get version, retry = 1(W37#5E330F7FF70)
2019.09.19 17:25:08 3: radino_433Mhz/init: MU;P0=-432;P1=849;P2=-861;P3=242;P4=-609;P5=602;P6=-243;D=01212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565;CP=5;R=0;O;
2019.09.19 17:25:19 3: radino_433Mhz/init: get version, retry = 2(W37#5E330F7FF70)
2019.09.19 17:25:19 3: radino_433Mhz/init: MU;P0=-432;P1=849;P2=-861;P3=242;P4=-609;P5=602;P6=-243;D=01212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565;CP=5;R=0;O;
2019.09.19 17:25:29 3: radino_433Mhz/init: get version, retry = 3(W37#5E330F7FF70)
2019.09.19 17:25:29 3: radino_433Mhz/init: MU;P0=-432;P1=849;P2=-861;P3=242;P4=-609;P5=602;P6=-243;D=01212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565656565634565656341212121234563456565656343434565634345656343434345656565634565656565656565;CP=5;R=0;O;
2019.09.19 17:25:29 2: radino_433Mhz/init retry count reached. Closed
2019.09.19 17:25:29 2: radino_433Mhz closed

2019.09.19 22:00:54 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.19 22:00:54 3: radino_433Mhz/init: get version, retry = 0(P7#B40067F19)
2019.09.19 22:00:54 3: radino_433Mhz/init: MS;P0=516;P1=-928;P3=-1958;P4=-3952;D=04030103030103010101010101010101010103030101030303030303030101010303010103;CP=0;SP=4;R=218;
2019.09.19 22:01:00 1: /dev/ttyACM1 disconnected, waiting to reappear (radino_433Mhz)
[... cut ...]
2019.09.19 22:01:01 1: /dev/ttyACM1 reappeared (radino_433Mhz)
2019.09.19 22:01:02 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.19 22:01:03 3: radino_433Mhz/init: get version, retry = 0(P7#B40067F19)
2019.09.19 22:01:03 3: radino_433Mhz/init: MS;P0=516;P1=-928;P3=-1958;P4=-3952;D=04030103030103010101010101010101010103030101030303030303030101010303010103;CP=0;SP=4;R=218;
2019.09.19 22:01:03 2: radino_433Mhz: initialized. v3.4.1_dev_09.09
2019.09.19 22:01:03 3: radino_433Mhz/init: enable receiver (XE)
```

HomeAutoUser, 2019-09-20:
@elektron-bbs, zur Info
der Reset scheint zu funktionieren beim radino mit deiner Version. Hier der Beweis, heute 12 Uhr "ging er Mittag machen" und dann ging der Empfang weiter ohne das sich die Schnittstelle geschlossen hatte. Aktuell läuft dieser auch noch normal.

```
2019.09.20 12:00:42 3: Opening radino_433Mhz device /dev/ttyACM1
2019.09.20 12:00:42 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.20 12:00:42 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.20 12:00:42 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.20 12:00:42 3: radino_433Mhz device opened
2019.09.20 12:00:44 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.20 12:00:44 3: radino_433Mhz/init: get version, retry = 0(W37#5E310F7DF70)
2019.09.20 12:00:44 3: radino_433Mhz/init: MU;P0=-850;P1=861;P2=236;P3=-603;P4=615;P5=-244;D=01010102345234545454523232345452323234523232323454545452345454545452345454545452345454523101010102345234545454523232345452323234523232323454545452345454545452345454545452345454523101010102345234545454523232345452323234523232323454545452345454545452345454;CP=2;R=6;O;
2019.09.20 12:00:50 1: /dev/ttyACM1 disconnected, waiting to reappear (radino_433Mhz)
2019.09.20 12:00:50 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.20 12:00:50 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.20 12:00:50 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.20 12:00:50 1: /dev/ttyACM1 reappeared (radino_433Mhz)
2019.09.20 12:00:52 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.20 12:00:52 3: radino_433Mhz/init: get version, retry = 0(W37#5E310F7DF70)
2019.09.20 12:00:52 3: radino_433Mhz/init: MU;P0=-850;P1=861;P2=236;P3=-603;P4=615;P5=-244;D=01010102345234545454523232345452323234523232323454545452345454545452345454545452345454523101010102345234545454523232345452323234523232323454545452345454545452345454545452345454523101010102345234545454523232345452323234523232323454545452345454545452345454;CP=2;R=6;O;
2019.09.20 12:00:52 2: radino_433Mhz: initialized. v3.4.1_dev_09.09
2019.09.20 12:00:52 3: radino_433Mhz/init: enable receiver (XE)
2019.09.20 13:26:01 3: Timer: time difference too large! interval=59, Sekunde=01
2019.09.20 13:45:01 3: Timer: time difference too large! interval=59, Sekunde=01
2019.09.20 14:04:01 3: Timer: time difference too large! interval=59, Sekunde=01
2019.09.20 15:05:04 3: radino_433Mhz: SD_WS_33 Parse msg W33#11E3F182446 - ERROR check 13 != 8
```

HomeAutoUser, 2019-09-20:
@elektron-bbs, zwecks flashen,
er versucht das ganze durchzuführen aber ich habe hinterher immer noch die selbe Version drauf obwohl ich eine andere zum flashen auswählte. Laut der optischen Kontrolle, flasht er auch aber die Version bleibt die gleiche.

```
2019.09.20 21:06:40 3: SIGNALduino_Set flash 3.3.1-RC10 try to fetch github assets for tag 3.3.1-RC10
2019.09.20 21:06:40 3: SIGNALduino_Set flash 3.3.1-RC10 try to fetch release https://api.github.com/repos/RFD-FHEM/SIGNALDuino/releases/tags/3.3.1-RC10
2019.09.20 21:06:47 3: radino_433Mhz: error while requesting https://github-production-release-asset-2e65be.s3.amazonaws.com/27504581/91ce2080-0b13-11e9-9555-2f2d5354eba0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190920%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190920T190641Z&X-Amz-Expires=300&X-Amz-Signature=b874bc44b69922b1f78ce653a84c797bdb06394a99929caf7a19f7a3beecaf4e&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DSIGNALDuino_radinocc1101R3.3.1-RC10.hex&response-content-type=application%2Foctet-stream - read from https://github-production-release-asset-2e65be.s3.amazonaws.com:443 timed out
2019.09.20 21:07:35 1: radino_433Mhz: found availableFirmware for ESP_1M,ESP32,nano328,nanoCC1101,miniculCC1101,promini,radinoCC1101
2019.09.20 21:07:50 3: SIGNALduino_Set flash 3.3.1-RC10 try to fetch github assets for tag 3.3.1-RC10
2019.09.20 21:07:50 3: SIGNALduino_Set flash 3.3.1-RC10 try to fetch release https://api.github.com/repos/RFD-FHEM/SIGNALDuino/releases/tags/3.3.1-RC10
2019.09.20 21:07:52 3: url https://github-production-release-asset-2e65be.s3.amazonaws.com/27504581/91ce2080-0b13-11e9-9555-2f2d5354eba0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190920%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190920T190751Z&X-Amz-Expires=300&X-Amz-Signature=694f8514306b18bf89b747a9a5f4083afbf651bc3c3454b63688b4c42d322e8c&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DSIGNALDuino_radinocc1101R3.3.1-RC10.hex&response-content-type=application%2Foctet-stream returned: 77974 bytes Data
2019.09.20 21:07:52 3: radino_433Mhz: Downloaded SIGNALDuino_radinocc1101R3.3.1-RC10.hex firmware from github-production-release-asset-2e65be.s3.amazonaws.com
2019.09.20 21:07:52 3: radino_433Mhz: filename FHEM/firmware/SIGNALDuino_radinocc1101R3.3.1-RC10.hex provided, trying to flash
2019.09.20 21:08:00 3: Opening radino_433Mhz device /dev/ttyACM1
2019.09.20 21:08:00 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.20 21:08:00 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.20 21:08:00 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.20 21:08:00 3: radino_433Mhz device opened
2019.09.20 21:08:00 3: radino_433Mhz: Firmware update was succesfull
2019.09.20 21:08:01 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.20 21:08:02 3: radino_433Mhz/init: get version, retry = 0(P7#B40073F1C)
```

Wiederum der manuelle Reset via FHEM geht auch, da ich mit Uptime das ganze Kontrollieren kann. Nach Reset ist immer Uptime wieder rückgesetzt aber eben flashen geht nicht.

elektron-bbs, 2019-09-21:
Was zeigt die Datei "SIGNALduino-Flash.log"?
Ich vermisse in deinen Logs eine Ausgabe in der Art:
2019.09.21 09:29:58 3: sduinoACM/flash: forcing special reset for radinoCC1101 on /dev/ttyACM0
Diese hatte ich zusätzlich eingebaut.
Das Attribut "hardware" muss auf "radinoCC1101" gesetzt sein.

HomeAutoUser, 2019-09-21:
@elektron-bbs 
das Attribut war und ist gesetzt

>   hardware   radinoCC1101

Die Zeile

>2019.09.20 21:11:38 3: radino_433Mhz/reset: forcing special reset for radinoCC1101 on /dev/ttyACM1

kommt auch sporadisch, also ist deine Variante in der pm auch aktiv. Beim flashen kommt diese nicht.
```

2019.09.21 12:41:38 1: radino_433Mhz: UNDEFINED Sensor SD_WS07_TH detected, code SD_WS07_TH_805
2019.09.21 14:25:12 1: radino_433Mhz: UNDEFINED Sensor SD_WS07_TH detected, code SD_WS07_TH_B06
2019.09.21 14:32:00 3: radino_433Mhz/KeepAlive not ok, retry = 2 -> get ping
2019.09.21 14:33:00 3: radino_433Mhz/KeepAlive not ok, retry = 3 -> get ping
2019.09.21 14:34:00 3: radino_433Mhz/keepalive not ok, retry count reached. Reset
2019.09.21 14:34:00 3: radino_433Mhz/reset: radinoCC1101
2019.09.21 14:34:31 3: radino_433Mhz/reset: forcing special reset for radinoCC1101 on /dev/ttyACM1
2019.09.21 14:34:32 3: Opening radino_433Mhz device /dev/ttyACM1
2019.09.21 14:34:32 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.21 14:34:32 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.21 14:34:32 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.21 14:34:32 3: radino_433Mhz device opened
2019.09.21 14:34:34 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.21 14:34:34 3: radino_433Mhz/init: get version, retry = 0(W37#5E320F84C30)
2019.09.21 14:34:34 3: radino_433Mhz/init: MU;P0=-239;P1=620;P2=243;P3=-605;P4=851;P5=-858;D=01010102323232310232310102323232310102345454545231023101010102323231010232310232323232310101010102323232310232310102323232310102345454545231023101010102323231010232310232323232310101010102323232310232310102323232310102345454545231023101010102323231010232;CP=2;R=8;O;
2019.09.21 14:34:39 1: /dev/ttyACM1 disconnected, waiting to reappear (radino_433Mhz)
2019.09.21 14:34:44 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.21 14:34:44 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.21 14:34:44 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.21 14:34:44 1: /dev/ttyACM1 reappeared (radino_433Mhz)
2019.09.21 14:34:45 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.21 14:34:46 3: radino_433Mhz/init: get version, retry = 0(W37#5E320F84C30)
2019.09.21 14:34:46 3: radino_433Mhz/init: MU;P0=-239;P1=620;P2=243;P3=-605;P4=851;P5=-858;D=01010102323232310232310102323232310102345454545231023101010102323231010232310232323232310101010102323232310232310102323232310102345454545231023101010102323231010232310232323232310101010102323232310232310102323232310102345454545231023101010102323231010232;CP=2;R=8;O;
2019.09.21 14:34:46 2: radino_433Mhz: initialized. v3.4.1_dev_09.09
2019.09.21 14:34:46 3: radino_433Mhz/init: enable receiver (XE)
2019.09.21 15:26:51 1: radino_433Mhz: SD_WS_Parse UNDEFINED sensor SD_WS_85_THW detected, code SD_WS_85_THW_1
2019.09.21 15:28:51 1: radino_433Mhz: UNDEFINED Sensor SD_WS07_TH detected, code SD_WS07_TH_B46
2019.09.21 15:43:01 3: Timer: time difference too large! interval=59, Sekunde=01
```

Flashversuch:
```
2019.09.21 22:31:08 1: radino_433Mhz: found availableFirmware for ESP_1M,ESP32,nano328,nanoCC1101,miniculCC1101,promini,radinoCC1101
2019.09.21 22:31:23 3: SIGNALduino_Set flash nightly try to fetch github assets for tag nightly
2019.09.21 22:31:23 3: SIGNALduino_Set flash nightly try to fetch release https://api.github.com/repos/RFD-FHEM/SIGNALDuino/releases/tags/nightly
2019.09.21 22:31:25 3: url https://github-production-release-asset-2e65be.s3.amazonaws.com/27504581/0cb9e380-2804-11e9-8a2e-ff085f3dd39a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190921%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190921T203124Z&X-Amz-Expires=300&X-Amz-Signature=573f00364a88550aefa2e98e3e6cfadc65b93d895ce5b26db828d06daa8f2dfc&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DSIGNALDuino_radinocc1101nightly.hex&response-content-type=application%2Foctet-stream returned: 80032 bytes Data
2019.09.21 22:31:25 3: radino_433Mhz: Downloaded SIGNALDuino_radinocc1101nightly.hex firmware from github-production-release-asset-2e65be.s3.amazonaws.com
2019.09.21 22:31:25 3: radino_433Mhz: filename FHEM/firmware/SIGNALDuino_radinocc1101nightly.hex provided, trying to flash
2019.09.21 22:31:30 3: Opening radino_433Mhz device /dev/ttyACM1
2019.09.21 22:31:30 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2019.09.21 22:31:30 1: radino_433Mhz/define: /dev/ttyACM1@57600
2019.09.21 22:31:30 1: radino_433Mhz/init: /dev/ttyACM1@57600
2019.09.21 22:31:30 3: radino_433Mhz device opened
2019.09.21 22:31:30 3: radino_433Mhz: Firmware update was succesfull
2019.09.21 22:31:32 3: radino_433Mhz/init: disable receiver (XQ)
2019.09.21 22:31:33 3: radino_433Mhz/init: get version, retry = 0(W37#5E320F86070)
2019.09.21 22:31:33 3: radino_433Mhz/init: MU;P0=242;P1=-615;P2=620;P3=-237;P4=864;P5=-844;D=01010101232301010101010123232301454545450123012323232301010123230101230101010101232323232301010101232301010101010123232301454545450123012323232301010123230101230101010101232323232301010101232301010101010123232301454545450123012323232301010123230101230101;CP=0;R=6;O;
2019.09.21 22:31:36 2: radino_433Mhz: initialized. v3.4.1_dev_09.09
2019.09.21 22:31:36 3: radino_433Mhz/init: enable receiver (XE)

```

Flashlog:
```
avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

[... 28 lines cut ...]
[... cut ...]
Found programmer: Id = "Mu;��"; type = �
    Software Version = .M; Hardware Version = u.;
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: programmer did not respond to command: leave prog mode
avrdude: error: programmer did not respond to command: exit bootloader

avrdude done.  Thank you.
```

elektron-bbs, 2019-10-02:
@sidey79 
Ich habe die Variante mit stty jetzt mehrfach durchprobiert. Hier funktioniert es so einwandfrei:

DevIo_CloseDev($hash);
if ($hardware eq "radinoCC1101" && $^O eq 'linux') {
   $hash->{logMethod}->($name, 3, "$hash->{TYPE} $name/flash: forcing special reset for $hardware on $port");
   # Mit dem Linux-Kommando 'stty' die Port-Einstellungen setzen
   system("stty -F $port ospeed 1200 ispeed 1200");
   sleep(1);	# ohne funktioniert es nicht
}
$hash->{STATE} = "FIRMWARE UPDATE running";

Wenn du damit einverstanden bist, würde ich einen Pull request einleiten.

sidey79, 2019-10-02:
Ja mach das.
Ob der system call richtig ist, da bin ich mir nie ganz sicher.

Es funktioniert aber :)

elektron-bbs, 2019-11-03:
Reset sollte ja jetzt funktionieren - können wir das schließen?

closed 2020-01-13 by HomeAutoUser