RFD-FHEM/SIGNALDuino #124: xFSK - extension firmware

HomeAutoUser, 2020-04-04:
* For receiving and evaluating xFSK signals

Zusammenführend zu https://github.com/RFD-FHEM/RFFHEM/issues/814 werden hier alle Anpassungen der Firmware zusammengefasst, diskutiert und zusammengetragen.

Notwendig werden vorerst Anpassung aus
https://github.com/Ralf9/SIGNALDuino/commit/a4417812de764d58e55f523f689bbea34788e393
https://github.com/Ralf9/SIGNALDuino/commit/cec03d893fa067de176d80f539ab21694f20c237
https://github.com/Ralf9/SIGNALDuino/commit/9f11be77e7ff416cac5b2898c7943289e941b6b4
https://github.com/Ralf9/SIGNALDuino/commit/2d80ad85869c8acc1e8f4704a89e053f6172bed6
sein. @Ralf9 steht uns für Rückfragen zur Verfügung.

elektron-bbs, 2020-04-04:
Hast du diese Variante auch schon mit dem Radino getestet?
Ich frage deshalb, weil ich in Erinnerung habe, das dort nur noch ca. 200 Byte Flash frei waren. Ich fände es unschön, wenn zukünftig der Radino von Firmware-Updates ausgeschlossen wäre.

HomeAutoUser, 2020-04-04:
Es ist nicht das Ziel irgendwelche Firmware-Updates auszuschließen.

Die Integration soll schon so passieren, das beim Kompilieren dies „gefiltert“ wird. Da eh bestimmte HardwareKonstelationen mit FSK nichts anfangen können, so muss eh eine Hardwareabfrage / Eingrenzung erfolgen. Dort wird dann der Radio mit aufgenommen. Das würde ich nicht als Problem ansehen.

elektron-bbs, 2020-04-05:
Stimmt auch wieder - ohne CC110x funktioniert diese Variante sowieso nicht und dann wahrscheinlich auch weiterhin Radino ohne FSK-Unterstützung.

HomeAutoUser, 2020-07-04:
@elektron-bbs zur Info,
mir gelang die Umsetzung der Umschaltung der Modulation via FHEM.
Seit gestern lasse ich auch 3 von meinen 4 Empfängern mit der neuen FW laufen. Beim radino klappt auch die Umstellung problemlos.

elektron-bbs, 2020-07-04:
Hervorragend, empfängst du auch irgendwelche FSK-Sensoren, außer dem bereits bekannten Typ?

HomeAutoUser, 2020-07-04:
Beim nächtlichen Test beim Radino, welchen ich bewußt umstellte auf 868Mhz wurde in dem Mode nichts empfangen.
Nun werde ich mal den Test vollziehen mit dem "real" 868Mhz cc1101 beim Nano und alternativ werde ich bewußt mal den Sensor, dessen mode ich einstellte, etwas von der Reichweite heranziehen. Ich bin der Meinung im Forum kam schonmal eine solch Diskussion bzw Erfahrung auf. xFSK ist nicht so "weit" empfangbar oder anfälliger.

HomeAutoUser, 2020-07-04:
@elektron-bbs, zur Info und besseren Einschätzung :-)

Aufzeichnung ab 17:30

Fernbedienung | Zustand | letzter Empfang | RSSI | Anz.
-- | -- | -- | -- | --
Lacrosse_mode1_08 | T: 28.3
Lacrosse_mode2_04_unser | T: 26.4 H: 39 | 2020-07-04 18:00:58 | -64 | 55
Lacrosse_mode2_27 | T: 26.4 | 2020-07-04 18:00:42 | -102 | 17
Lacrosse_mode2_39 | T: 26.4 | 2020-07-04 17:55:17 | -100.5 | 17
Lacrosse_mode2_3A | T: 26.5 | 2020-07-04 18:00:47 | -104.5 | 5
Lacrosse_mode2_3F | T: 25.7 | 2020-07-04 17:52:20 | -100 | 27

Erkenntnis, Autocreate muss noch geprüft werden weil ich es manuell anlegte alles.

elektron-bbs, 2020-07-04:
Das mit dem Autocreate liegt sicher an dem "LaCrossePairForSec <sec> [ignore_battery]" - im Lacrosse-Modul.

sidey79, 2020-07-04:
Autocreate ist doch ein eigenes Modul.

Wenn wir die Daten auch im SIGNALduino-Tool haben, dann kann das automatisch mit geprüft werden

HomeAutoUser, 2020-07-04:
Also ich habe daheim nun auch die Umschaltung geprüft und diese funktioniert dort ebenso. :-)

>Das mit dem Autocreate liegt sicher an dem "LaCrossePairForSec sec [ignore_battery]" - im Lacrosse-Modul.

Vermutlich liegt dort noch ein anderer Fehler vor. Soeben kam gesetzten Timer keine Autoanlegung obwohl Autocreate aktiv ist.

>Wenn wir die Daten auch im SIGNALduino-Tool haben, dann kann das automatisch mit geprüft werden

Ja, das ist richtig aber es geht hier um die Funktion welche wir ja kurzzeitig aktivieren weil das Modul kein Autocreate unterstüzt wenn ich das richtig noch zusammen bekomme.

elektron-bbs, 2020-07-04:
Das Modul 36_LaCrosse.pm unterstützt schon autocreate:

`set LGW LaCrossePairForSec <sec> [ignore_battery]
`
`enable autocreate of new LaCrosse sensors for <sec> seconds. If ignore_battery is not given only sensorssending the 'new battery' flag will be created. 
`

Bei mir wurden die Sensoren auch automatisch angelegt. Ich schätze, wir müssten da ein Bit ändern, damit Sensoren auch angelegt werden ohne diesen Befehl.

HomeAutoUser, 2020-07-04:
Von der Ferne, denke ich ggf an die Umsetzung von uns im Signalduino Modul. Dort wird vermutlich nicht der Timer mit allen Funktionen gestartet. Dort möchte ich demnächst ansetzen. Nach einem manuellen Anlegen kommen ja die Nachrichten durch.

sidey79, 2020-07-04:
Wo muss denn ein Timer gestartet werden?

Autocreate wird doch vom logischen Modul gesteuert. Wir liefern doch "nur" die Daten an das logische Modul.

elektron-bbs, 2020-07-05:
Diesen Befehl setze ich z.B. im LaCrosseGateway:

`set LGW LaCrossePairForSec <sec> [ignore_battery]
`

In der 36_LaCrosseGateway.pm wird daraufhin ein Timer gestartet und ein Wert in den Hash geschrieben:

```
  elsif ($cmd eq "LaCrossePairForSec") {
    my @args = split(' ', $arg);

    return "Usage: set $name LaCrossePairForSec <seconds_active> [ignore_battery]" if(!$arg || $args[0] !~ m/^\d+$/ || ($args[1] && $args[1] ne "ignore_battery") );
    $hash->{LaCrossePair} = $args[1]?2:1;
    InternalTimer(gettimeofday()+$args[0], "LaCrosseGateway_RemoveLaCrossePair", $hash, 0);
  }
```

In der 36_LaCrosse.pm wird autocreate nur angeworfen, wenn battery_new vom Sensor kommt oder LaCrossePair im Hash auf 2 gesetzt wuurde:

`    return "UNDEFINED LaCrosse_$rname LaCrosse $raddr" if( $battery_new || $hash->{LaCrossePair} == 2 );
`

HomeAutoUser, 2020-07-05:
> set LGW LaCrossePairForSec <sec> [ignore_battery] 

Dieses haben wir ja integriert aber die Umsetzung ist noch fehlerhaft.

```
2020.07.05 14:28:10 4: nano_868Mhz: Set_LaCrossePairForSec, LaCrosse autocreate active for 60 seconds
2020.07.05 14:28:18 4: nano_868Mhz: Read, msg: MN;D=9A46606AB0F99DF8459BDED9BDFCDAE5FBDECFA79297FFEAF8D4B06CF763DB10;R=246;
2020.07.05 14:28:18 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2020.07.05 14:28:18 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 100 dmsg=OK 9 41 1 4 236 106 RSSI = -79
2020.07.05 14:28:18 4: LaCrosse: Unknown device 29, please define it
2020.07.05 14:28:18 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 103 -> Lacrosse mode 2 with match (?^:^9)
2020.07.05 14:28:18 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 103 dmsg=OK 9 41 1 4 236 106 RSSI = -79
2020.07.05 14:28:18 4: nano_868Mhz: Dispatch, OK 9 41 1 4 236 106, Dropped due to short time or equal msg
2020.07.05 14:28:28 4: nano_868Mhz: Read, msg: MN;D=9A46606AB0C99ED91945086FEFFAA9BFDA85B472BCF89604E6BCB50E763DEFFE;R=247;
2020.07.05 14:28:28 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2020.07.05 14:28:28 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 100 dmsg=OK 9 41 1 4 236 106 RSSI = -78.5
2020.07.05 14:28:28 4: LaCrosse: Unknown device 29, please define it
2020.07.05 14:28:28 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 103 -> Lacrosse mode 2 with match (?^:^9)
2020.07.05 14:28:28 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 103 dmsg=OK 9 41 1 4 236 106 RSSI = -78.5
2020.07.05 14:28:28 4: nano_868Mhz: Dispatch, OK 9 41 1 4 236 106, Dropped due to short time or equal msg
2020.07.05 14:28:35 4: nano_868Mhz: KeepAlive, ok, retry = 0
2020.07.05 14:28:48 4: nano_868Mhz: Read, msg: MN;D=9A46606AB0FF76FACC6589C7F33CCA6FBEF6BF2E6EFF32EA7FCBFCD0D7FDE3F1;R=246;
2020.07.05 14:28:48 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2020.07.05 14:28:48 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 100 dmsg=OK 9 41 1 4 236 106 RSSI = -79
2020.07.05 14:28:48 4: LaCrosse: Unknown device 29, please define it
2020.07.05 14:28:48 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 103 -> Lacrosse mode 2 with match (?^:^9)
2020.07.05 14:28:48 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 103 dmsg=OK 9 41 1 4 236 106 RSSI = -79
2020.07.05 14:28:48 4: nano_868Mhz: Dispatch, OK 9 41 1 4 236 106, Dropped due to short time or equal msg
2020.07.05 14:28:58 4: nano_868Mhz: Read, msg: MN;D=9A46606AB037EDF7CE8DB92C7ED6DABC887E6FB6E6DCFBAFD6DBFA4A155FF6CC;R=245;
2020.07.05 14:28:58 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2020.07.05 14:28:58 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 100 dmsg=OK 9 41 1 4 236 106 RSSI = -79.5
2020.07.05 14:28:58 4: LaCrosse: Unknown device 29, please define it
2020.07.05 14:28:58 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 103 -> Lacrosse mode 2 with match (?^:^9)
2020.07.05 14:28:58 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 103 dmsg=OK 9 41 1 4 236 106 RSSI = -79.5
2020.07.05 14:28:58 4: nano_868Mhz: Dispatch, OK 9 41 1 4 236 106, Dropped due to short time or equal msg
2020.07.05 14:29:08 4: nano_868Mhz: Read, msg: MN;D=9A46606AB045C6D515927F47EFCCEE6ECD71DBA62D92572D667DFFF7BE56B93F;R=246;
2020.07.05 14:29:08 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2020.07.05 14:29:08 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 100 dmsg=OK 9 41 1 4 236 106 RSSI = -79
2020.07.05 14:29:08 4: LaCrosse: Unknown device 29, please define it
2020.07.05 14:29:08 4: nano_868Mhz: Parse_MN, Found 2-FSK Protocol id 103 -> Lacrosse mode 2 with match (?^:^9)
2020.07.05 14:29:08 4: nano_868Mhz: Parse_MN, Decoded matched MN Protocol id 103 dmsg=OK 9 41 1 4 236 106 RSSI = -79
2020.07.05 14:29:08 4: nano_868Mhz: Dispatch, OK 9 41 1 4 236 106, Dropped due to short time or equal msg
2020.07.05 14:29:10 4: nano_868Mhz: Set_LaCrossePairForSec, time expired, LaCrosse autocreate deactivate
```

Im Device nano_868Mhz existiert auch im hast LaCrossePair 1.

HomeAutoUser, 2020-07-05:
Fehler gefunden.
`	$hash->{LaCrossePair} = 1;  # LaCrosse autoCreateState: 0 = autoreate not defined | 1 = autocreate defined | 2 = autocreate active`

das muss eine 2 werden. Da haben wir beim integrieren nicht richtig aufgepasst.

`	$hash->{LaCrossePair} = 2;  # LaCrosse autoCreateState: 0 = autoreate not defined | 1 = autocreate defined | 2 = autocreate active`

funktioniert.

Das Modul regiert ja nur auf eine 2
`	return "UNDEFINED LaCrosse_$rname LaCrosse $raddr" if( $battery_new || $hash->{LaCrossePair} == 2 );`

elektron-bbs, 2020-07-05:
Ach, wie gut, das ich gegen die Übernahme von dev-3.5 in den Master war :-)

sidey79, 2020-07-05:
Wieso?

elektron-bbs, 2020-07-05:
Weil eben der nächste Bug entdeckt wurde.

sidey79, 2020-07-05:
Ist was abgestürzt oder ähnliches?
Ich verstehe gerade nicht, was die Diskussion mit der Firmware angeht.

Was dieses seltsame Internal angeht. Da haben wir schon diskutiert, dass es nicht in das IODev gehört Autocreate zu steuern.

elektron-bbs, 2020-07-05:
Du hast Recht, das es nicht zur Firmware gehört.
Wenn wir das Lacrosse-Modul verwenden wollen, müssen wir uns schon anpassen. Lies bitte nochmal meinen Beitrag: https://github.com/RFD-FHEM/SIGNALDuino/issues/124#issuecomment-653866425

HomeAutoUser, 2020-07-05:
Schluss mit den kleinen Sticheleien ;-)

Es wäre schön wenn sich nicht zu viel aufstaut an PR’s ist meine Meinung. Es geht auch darum um den Durchblick zu behalten. Desto mehr wir einbringen, an mehr Stellen diskutieren / ändern wir mit dem Konsens, keiner sieht mehr so richtig durch und es bleibt länger liegen :-D

HomeAutoUser, 2020-08-04:
@sidey79,
sehe ich das richtig, das wir hier https://github.com/RFD-FHEM/SIGNALDuino/releases
nur die Dateien bereit stellen ohne die Kompilfunktion Debug?

Ich frage aus folgendem Grund.
Derzeit bieten wir die Möglichkeit an via PlatformIO Ini Datei, sämtliche uC´s mit den Debug Funktionen.
Durch die Erweiterung des Codes kommen nun wenige uC´s bei eingeschalteter Debug Option an die Grenze des verwendbaren.
--> Also geht es nicht zu kompilieren weil der Speicher zu ca. 2-6% überschritten wird. (normal ohne Debug funktioniert es)

Bisher habe ich diese Information festgehalten in der `compile_config.h` Datei. Ich denke, diese auch aus der PaltformIO Ini Datei auszukommentieren um nicht die Option anzubieten wenn diese eh fehl schlagen würde. Wie denkt Ihr darüber?

Eine andere Möglichkeit ist vielleicht, zusätzlich die xFSK Möglichkeit als `#define` Variable zu erstellen. Wie undurchsichtig oder Übersichtlich dies dann wird, habe ich mir noch nicht angesehen.

Ich leite vorerst den Test der Firmware auf meinen Hardwaregeräten ein weil ich Erfolge erzielte. *smile*

HomeAutoUser, 2020-09-05:
@sidey79 
ich nehme nun den ganzen Code unter die Lupe und sehe einige Leichen vermutlich oder Fehler welche untergingen.

Grund dafür, das ich versuche alle Variablen zu setzen beim Compilieren ist um das Ergebnis zu vergleichen.
Versuche bitte die Firmware mit dem ESP8266 zu compilieren und setze hierfür

1) compile_config.h
#define OTHER_BOARD_WITH_CC1101  1                             // boards with CC1101 (example: ESP8266, ESP32, Maple Mini ...)
#define DEBUG

2) signalDecoder.h
#define DEBUG 1
#define DEBUGDECODE 1

Es folgt ein Fehler 
```
exit status 1
invalid conversion from '__FlashStringHelper*' to 'uint8_t {aka unsigned char}' [-fpermissive]
```

Kann es sein, das beim ESP der __FlashStringHelper anders definiert wird?
Hier https://github.com/RFD-FHEM/SIGNALDuino/blob/master/src/_micro-api/libraries/output/src/output.h#L43:L45 wird dieser nur für nicht ESP definiert.

Bitte mal drüber schauen.
Ziel sollte es sein, das es fehlerfrei zu combilieren geht, sonst bringen wir immer mehr Leichen in den Code und dieser ist unnötig aufgequollen.

HomeAutoUser, 2020-09-05:
Der Fehler mit 
```
exit status 1
invalid conversion from '__FlashStringHelper*' to 'uint8_t {aka unsigned char}' [-fpermissive]
```
wurde von mir gefunden und es war nur eine Zeile wo ein falscher Parameter mit vorhanden war.

Ich werde mir vermutlich mal 2 zusätzliche DEFs (DEBUGDETECT / DEBUGDECODE) in Platform IO einbinden um auch diese Dinge ständig im Auge zu behalten, das dort kein Fehler vorkommt.

sidey79, 2020-09-05:
Ja, ich hatte einiger dieser Flash String Probleme beseitigt, aber damals nicht mit den erweiterten Debug Optionen geprüft.

HomeAutoUser, 2020-09-05:
Gibt es eine Möglichkeit wie ich die Verarbeitung FHEM <--> ESP8266 Debuggen kann ähnlich wie beim kabelgebundenem Nano?

sidey79, 2020-09-06:
Meinst Du die Seriellen Ausgaben auf dem Nano?

Kommen die nicht wie beim Nano auf dem TCP Port 23 mit?

HomeAutoUser, 2020-09-06:
Daran dachte ich noch nicht, es auf IP + Port 23 zu versuchen. Da sollte ich ja auch die Befehle setzen können. Ich versuche es mal.

HomeAutoUser, 2021-05-26:
Die Anpassung für "xFSK - extension firmware" ist in der Firmware eingebunden und so schließe ich den Faden.

closed 2021-05-26 by HomeAutoUser