RFD-FHEM/SIGNALDuino #125: Bekomme SignalDuino nicht auf NodeMCU2 mit CC1101 compiliert

Dattel, 2020-05-17:
Hi,
habe vergeblich versucht, SIgnalDuino für mein NodeMCU2 mit CC1101 zu kompilieren.
Bisher nutze ich erfolgreich die veraltete SignalESP-Version auf dem selben Board.
Vielleicht kann mir da jemand Hilfestellung geben. Ich nurte VS-Code mit dem Platformio-Plugin
Folgendes habe ich gemacht:

in der platformio.ini
```
[platformio]
default_envs = esp8266cc1101@latest-debug
```
kompiliert und hochgeladen bekomme ich das ganze, aber es scheint, als wenn irgendwas mit den #Defines nicht stimmt denn:

In FHEM meldet sich mein Board mit dem Namen: "SIGNALDuino" anstelle von "SIGNALDESP", und der CC1101 Chip wird auch nicht initialisiert. Wenn ich in der compile_config.h explizit #define CMP_CC1101 setze, dann bekomme ich Compilierfehler (sämtliche CC1101::Funktionen werden hier als undefined-Fehler ausgeschmissen)... Also habe ich mal probiert, und habe #define CMP_CC1101 ganz oben in der cc1101.cpp gesetzt. Dann kann ich erfolgreich kompilieren und der Chip wird auch initialisiert. Aber das ist ja nicht Sinn der Sache, oder? Was mache ich falsch?

Gruß
Dattel

sidey79, 2020-05-17:
Hi,

Hast Du den Branch https://github.com/RFD-FHEM/SIGNALDuino/tree/dev-r3.4_plattformIO?files=1 verwendet?

Dattel, 2020-05-17:
gute Frage.... ich weiß, dass ich den bräuchte:
hab den main ausgecheckt 
`git clone https://github.com/RFD-FHEM/SIGNALDuino.git`

und dann im vsCode auf die '3.4.0-dev+20200413' geswitched.

Dattel, 2020-05-17:
ahh... bin ich blind.. ich probiere es gerade mit der korrekten branch

Dattel, 2020-05-17:
Leider nein... 
ich bekomme esp8266cc1101@latest sauber kompiliert und hochgeladen, aber der CC1101 scheint nicht initialisiert zu werden.

Pack ich dann in der compile.config.h noch `#define OTHER_BOARD_WITH_CC1101  1` rein, dann kommen die Fehler beim compilieren:
```....
undefined reference to `cc1101::regCheck()'
...
```
Wenn ich jetzt in der cc1101.cpp oben `#define CMP_CC1101` reinpacke, kann ich wieder kompilieren und auch sauber uploaden... 

Wundert mich, da in der compile_config.h ja explizit drin steht:

```
#ifdef OTHER_BOARD_WITH_CC1101
#define CMP_CC1101     
#endif
```
Außerdem hätte das Ganze ja ohne jegliche Änderungen funktionieren müssen, da in der platformio.ini `build_flags=-D DOTHER_BOARD_WITH_CC1101` drine sind

sidey79, 2020-05-17:
Die Angabe in der compile_config reicht leider nicht aus um den Compiler richtig zu instruieren.

Dafür ist die Option
build_flags=-D

Enthalten. Habe ich richtig verstanden, dass damit esp8266cc1101@latest richtig compiliert?

Wenn Du schreibst, dass der cc1101nicht richtig initialisiert wird. Wie macht sich das für dich bemerkbar?

Dattel, 2020-05-17:
also... nochmal mein Vorgehen:

- frisch aus dem repository gecloned
- `[platformio] default_envs = esp8266cc1101@latest`
- Compiliert und geuploaded, ins WLAN genommen -> in FHEM sind die ganzen CC1101 Readings ohne Funktion (cc1101_patable,cc1101_config_ext, cc1101_config), auch fehlt im VersionString dann der Vermerk (chip CC1101)

- Wenn ich dann in der compile_config.h explizit nochmal den Prepozessorbefehl `#define OTHER_BOARD_WITH_CC1101  1` einkommentiere, dann bekomme ich Compilerfehler über nicht deklarierte cc1101::funktionen
- diese bekomme ich weg, wenn ich oben in der cc1101.cpp explizit nochmal `#define CMP_CC1101`
reinhaue.
-wenn ich dann dieses Kompilat dann auf den ESP uploade, dann gehen auch die CC1101 Readings in FHEM sauber und ich empfange auch was.

Mir scheint, als wenn die Präprozessordefinitionen irgendwie in der ??falschen Reihenfolge?? beim Linken abgearbeitet werden bzw das die aus der platformio.ini auch nicht wirklich ziehen bei mir...

Dattel, 2020-05-17:
okay, mit den folgenden BuildFlags in der platformio.ini geht's

`build_flags=-D OTHER_BOARD_WITH_CC1101 -D CMP_CC1101`

keine Ahnung, warum es ohne nicht funktioniert

Dattel, 2020-06-06:
Feedback: 
das Ganze läuft im Übrigen deutlich stabiler als der ehemalige Fork direkt für den ESP. (warum auch immer - ist ja die selbe Code-Basis)
Ich betreibe das Ganze auf einen LOLIN ESP8266 und musste vorher einmal die Woche einen Reboot fahren, da sich der SignalESP aufgehängt hat..... Also nochmal Danke für's Zusammenführen..

BTW: vielleicht kannste ja nochmal für die nächsten Branch die BuildFlags in der Platform.io korrigieren

Gruß und Close...

closed 2020-06-06 by Dattel