RFD-FHEM/SIGNALDuino #126: [Feature Request] Support for RFM12/RFM69 module

eku, 2020-05-18:
The RFM12/69 modules can be configured in OOK mode and thus operated like a stupid receiver module on the Arduino Nano. An example could be the reception of FS20 signals in Ethersex (https://github.com/ethersex/ethersex/blob/master/hardware/radio/rfm12/rfm12_fs20.c). With a little adjustment, you could then perhaps also operate the popular JeeLinks v3/v3c with SIGNALduino.

sidey79, 2020-06-10:
Ist there a benefit yousing the rfm module instead of a cc1101?

I am asking because, adding Support for additional module rises complexity and needs some time.

eku, 2020-06-10:
@sidey79  yes, there is. You could use [JeeLink](https://jeelabs.net/projects/hardware/wiki/jeelink) with SIGNALduino.