RFD-FHEM/SIGNALDuino #127: SIGNALDuino crashes after 1-2 weeks

fredlcore, 2020-06-07:
I'm running SIGNALDuino with this firmware:
V 3.4.0-dev SIGNALduino cc1101 (chip CC110 unknown) - compiled at Dec 4 2019 22:02:15
I'm using the development version in order to access my EFTH-800 temperature sensors. This works great, but after 1-2 weeks the Nano seems to hang, neither issueing a "set reset" from FHEM nor opening a terminal connection and issueing a hangup brings the Nano back.
Once I unplug and reconnect the USB cable, everything is fine - again for a few weeks.

Is this a known problem or anything I can do to assist finding the bug?

sidey79, 2020-06-10:
Currently there ist no known bug which causes this.

In the past, there was a bug in the firmware which caused a reboot.
Most times, problems like this are caused by hardware problems like the arduino itself or the usb cable.

Die you have this problem also with Version 3.3.1 ?

fredlcore, 2020-06-10:
Hm, it would be difficult to debug because my sensors are not supported in 3.3.1 and I would have to run it for a couple of weeks to be certain that the error does not occur here.
The Nano is connected directly to the USB port, so there is no cable involved. As for problems with the Arduino itself, I wonder what kind of faulty hardware that should be if it runs fine for weeks. Based on the project that I've developed on an Arduino Mega (bsb-lan.de) which runs on several hundred devices, something like this looks like a software issue to me (either Firmware or FHEM module).

When I connect to /dev/ttyUSB0 via minicom and issue a hangup, there are regularly coming in binary messages on the serial port. Here is a log of about five minutes:
```
02 4d 75 3b b0 f2 80 3b 91 ee 80 3b b2 f0 81 3b 93 ea 81 3b a4 8c 93 3b 95 dc 82
 3b b6 da 82 3b 44 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 0
3 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 45 65 
65 65 61 21 21 23 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03
 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 3b 43 3
1 3b 52 31 42 3b 03 0a 3b 53 25 58 3b 0d 0a 02 4d 75 3b b0 e1 80 3b 91 f7 81 3b 
92 fb 80 3b b3 e4 81 3b a4 90 93 3b 95 df 82 3b b6 d9 82 3b 44 01 02 32 32 31 01
 02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 0
2 31 01 01 01 02 32 31 02 32 32 45 65 65 65 62 32 32 32 32 31 01 02 32 32 31 01 
02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02
 31 01 01 01 02 32 31 02 32 32 3b 43 32 3b 52 31 32 3b 03 0a 02 4d 75 3b b0 f4 8
0 3b 91 f0 80 3b b2 ee 81 3b 93 e3 81 3b b4 fc 92 3b 95 ec 82 3b b6 d5 82 3b 44 
01 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21
 21 21 21 23 03 03 01 23 01 23 01 23 03 03 01 23 01 23 01 45 65 65 65 61 21 23 0
1 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 
21 21 21 23 03 03 01 23 01 23 01 23 03 03 01 23 01 23 01 3b 43 31 3b 52 32 41 3b
 03 0a 02 4d 75 3b b0 ee 80 3b 91 f6 80 3b b2 e1 81 3b 93 f2 81 3b a4 94 93 3b 9
5 e0 82 3b b6 d8 82 3b 44 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 
23 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21
 45 65 65 65 61 21 21 23 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 2
3 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 
3b 43 31 3b 52 31 41 3b 03 0a 02 4d 75 3b b0 f0 80 3b 91 f7 81 3b 92 f1 80 3b b3
 e0 81 3b a4 80 93 3b 95 e1 82 3b b6 d8 82 3b 44 01 02 32 32 31 01 02 32 32 32 3
2 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 
02 32 31 02 32 32 45 65 65 65 62 32 32 32 32 31 01 02 32 32 31 01 02 32 32 32 32
 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 0
2 32 31 02 32 32 3b 43 32 3b 52 31 33 3b 03 0a 02 4d 75 3b b0 f3 80 3b 91 f7 80 
3b b2 df 81 3b 93 ee 81 3b a4 90 93 3b 95 ea 82 3b b6 d4 82 3b 44 01 21 21 23 01
 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 0
3 03 01 23 01 23 01 23 03 03 01 23 01 23 01 45 65 65 65 61 21 23 01 21 21 23 01 
21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03
 03 01 23 01 23 01 23 03 03 01 23 01 23 01 3b 43 31 3b 52 32 42 3b 03 0a 02 4d 7
5 3b b0 fd 80 3b 91 f2 80 3b b2 e8 81 3b 93 e3 81 3b a4 88 93 3b 95 ea 82 3b b6 
d7 82 3b 44 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01
 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 45 65 65 65 6
1 21 21 23 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 
23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 3b 43 31 3b 52
 31 42 3b 03 0a 02 4d 75 3b b0 ee 80 3b 91 e8 81 3b 92 f1 80 3b b3 e5 81 3b a4 a
0 93 3b 95 da 82 3b b6 dd 82 3b 44 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 
02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32
 32 45 65 65 65 62 32 32 32 32 31 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 0
2 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 
32 3b 43 32 3b 52 31 32 3b 03 0a 02 4d 75 3b b0 ed 80 3b 81 81 81 3b b2 e6 81 3b
 93 e8 81 3b b4 fc 92 3b 95 ea 82 3b b6 cf 82 3b 44 01 21 21 23 01 21 23 01 23 0
1 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 
23 01 23 03 03 01 23 01 23 01 45 65 65 65 61 21 23 01 21 21 23 01 21 23 01 23 01
 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 2
3 01 23 03 03 01 23 01 23 01 3b 43 31 3b 52 32 41 3b 03 0a 02 4d 75 3b b0 e8 80 
3b 91 f5 80 3b b2 e4 81 3b 93 ef 81 3b a4 9c 93 3b 95 e3 82 3b b6 d8 82 3b 44 01
 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 2
1 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 45 65 65 65 61 21 21 23 01 
21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 21
 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 3b 43 31 3b 52 31 43 3b 03 0
a 02 4d 75 3b b0 fb 80 3b 91 e2 81 3b 92 ee 80 3b b3 f1 81 3b a4 80 93 3b 95 e0 
82 3b b6 d9 82 3b 44 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32
 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 45 65 65 6
5 62 32 32 32 32 31 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32 
32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 3b 43 32 3b
[... cut ...]
;S%X; 
```

Rebooting the NAS (which includes FHEM) does not help. Only when I unplug and reinsert the Nano, FHEM immediately detects the status as "open". At the same time, I get this output
```
;S%X;
V 3.4.0-dev SIGNALduino cc1101 (chip CC110 unknown) - compiled at Dec  4 2019 25
```

So in ASCII, the fist line is similar to the output of the "hung" Nano. So it seems to me that resetting the Nano only gets it to a certain point where it hangs and only physically removing power can solve this issue.

Ralf9, 2020-06-11:
Was ergibt ein "get sduino ccreg 31"?

Evtl hast Du eine fehlerhafte Hardware, z.B. ein nicht orginaler FTDI oder den Test Pin Fehler
https://forum.fhem.de/index.php/topic,101785.msg952934.html#msg952934

fredlcore, 2020-06-11:
`unknown Register 31, please choose a valid cc1101 register`

Ralf9, 2020-06-11:
bitte versuch mal ein "set sduino raw C31"
Damit Du die Rückgabe im log siehst, musst Du vorher das verbose auf 4 setzen.

Du kannst auch in einem seriellen Terminal den Befehl C31 eingeben

fredlcore, 2020-06-11:
> Evtl hast Du eine fehlerhafte Hardware, z.B. ein nicht orginaler FTDI oder den Test Pin Fehler

Der Rechner (bzw. die NAS) erkennt den Nano ja durchgehend. Ich komme, wie oben zu sehen, ja auch nach dem "Absturz" weiterhin auf die serielle Konsole und bekomme dort auch (für mich nicht nachvollziehbare, aber zumindest repetitive) Daten ausgegeben. Ein problematischer FTDI wird es von daher wohl nicht sein, und der Test Pin Fehler scheint, wenn ich die verlinkten Artikel jetzt richtig überflogen habe, geht es da ja um USB-Fehlermeldungen in den Logs oder ein nicht erkanntes Gerät. Im Gegenteil löst ja ein erneutes Anstecken sofort das Problem.

fredlcore, 2020-06-11:
> bitte versuch mal ein "set sduino raw C31"

```
2020.06.12 01:03:02 3: SignalDuino: Attr, setting Verbose to: 4
2020.06.12 01:03:03 4: SignalDuino: Read, msg READredu: MU;P0=-222;P1=262;P2=-485;P3=500;P4=-4864;P5=745;P6=-714;D=012121230121230123012121212123012123030123012301230121212121230303012123012121212121230123014565656561212301212123012123012301212121212301212303012301230123012121212123030301212301212121212123012301;CP=1;R=43;
2020.06.12 01:03:03 4: SignalDuino: Parse_MU, Fingerprint for MU Protocol id 27 -> EFTH-800 matches, trying to demodulate
2020.06.12 01:03:03 4: SignalDuino: Parse_MU, Decoded matched MU Protocol id 27 dmsg W27#22504D507205 length 48 dispatch(1/4) RSSI = -52.5
2020.06.12 01:03:03 4: SignalDuino: SD_WS_Parse protocol 27, rawData 22504D507205
2020.06.12 01:03:03 4: SignalDuino: SD_WS_27 Parse msg 22504D507205, CRC 05
2020.06.12 01:03:03 4: SignalDuino: SD_WS_Parse decoded protocol-id 27 (EFTH-800), sensor-id 250
2020.06.12 01:03:17 3: EthRelais: Read callback: Error: 192.168.1.50: No route to host
2020.06.12 01:03:17 4: SignalDuino: Read, msg READredu: MU;P0=-12562;P1=504;P2=745;P3=-717;P4=249;P5=-468;P6=-239;P7=-4868;D=0102323232345454545451616454545161645454545451645451645164545161616454545451645454545164545451616451645451647232323234545454545161645454516164545454545164545164516454516161645454545164545454516454545161645164545164;CP=4;R=19;
2020.06.12 01:03:17 4: SignalDuino: Parse_MU, Fingerprint for MU Protocol id 27 -> EFTH-800 matches, trying to demodulate
2020.06.12 01:03:17 4: SignalDuino: Parse_MU, Decoded matched MU Protocol id 27 dmsg W27#06304A708469 length 48 dispatch(1/4) RSSI = -64.5
2020.06.12 01:03:17 4: SignalDuino: SD_WS_Parse protocol 27, rawData 06304A708469
2020.06.12 01:03:17 4: SignalDuino: SD_WS_27 Parse msg 06304A708469, CRC 69
2020.06.12 01:03:17 4: SignalDuino: SD_WS_Parse decoded protocol-id 27 (EFTH-800), sensor-id 630
2020.06.12 01:03:17 4: SignalDuino: Parse_MU, Decoded matched MU Protocol id 27 dmsg W27#06304A708469 length 48 dispatch(2/4) RSSI = -64.5
2020.06.12 01:03:17 4: SignalDuino: Dispatch, W27#06304A708469, Dropped due to short time or equal msg
2020.06.12 01:03:21 4: SignalDuino: Read, msg READredu: MU;P0=-255;P1=254;P2=-480;P3=481;P4=-4864;P5=731;P6=-742;D=0121212301212303030301212123012123030121212121230121212121230301212121230301212303012121214565656561212123012121230121230303030121212301212303012121212123012121212123030121212123030121230301212121;CP=1;R=30;
2020.06.12 01:03:21 4: SignalDuino: Parse_MU, Fingerprint for MU Protocol id 27 -> EFTH-800 matches, trying to demodulate
2020.06.12 01:03:21 4: SignalDuino: Parse_MU, Decoded matched MU Protocol id 27 dmsg W27#113C4C106198 length 48 dispatch(1/4) RSSI = -59
2020.06.12 01:03:21 4: SignalDuino: SD_WS_Parse protocol 27, rawData 113C4C106198
2020.06.12 01:03:21 4: SignalDuino: SD_WS_27 Parse msg 113C4C106198, CRC 98
2020.06.12 01:03:21 4: SignalDuino: SD_WS_Parse decoded protocol-id 27 (EFTH-800), sensor-id 13C
2020.06.12 01:03:24 4: SignalDuino: Set_raw, raw 31
2020.06.12 01:03:24 4: SignalDuino: HandleWriteQueue, called
2020.06.12 01:03:24 4: SignalDuino: SendFromQueue, called
2020.06.12 01:03:24 4: SignalDuino: Read, msg: Unsupported short command
2020.06.12 01:03:25 4: SignalDuino: HandleWriteQueue, called
2020.06.12 01:03:25 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.12 01:03:25 4: SignalDuino: KeepAlive, ok, retry = 0
```

Ralf9, 2020-06-11:
`set_raw, raw 31`
Das passt so nicht es muss heissen
set raw C31

fredlcore, 2020-06-12:
Sorry, hatte das "C" übersehen, hier der Output:
```
2020.06.12 08:45:25 4: SignalDuino: Set_raw, raw C31
2020.06.12 08:45:25 4: SignalDuino: HandleWriteQueue, called
2020.06.12 08:45:25 4: SignalDuino: SendFromQueue, called
2020.06.12 08:45:25 4: SignalDuino: Read, msg: C31 = 17
2020.06.12 08:45:25 4: SignalDuino: HandleWriteQueue, called
2020.06.12 08:45:25 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.12 08:45:26 4: SignalDuino: KeepAlive, ok, retry = 0
```

Ralf9, 2020-06-12:
Die cc1xxx Version 17 ist mir unbekannt, kann aber trotzem funktionieren.
Ich habe ein Modul mit der Version 18, das ohne Einschränkung funktioniert obwohl es kein cc1101 ist.

Kannst Du mal versuchen den cc1xxx chip auf dem 433 Mhz Modul zu fotografieren oder abzulesen, falls er nicht vom Schrumpfschlauch verdeckt ist.
Die Beschriftung auf dem 433 Mhz Modul ist auch interessant.

Ralf9, 2020-06-12:
Ich hab mal geschaut, der ccver nach müsste es der CC110L sein
https://www.ti.com/lit/ds/symlink/cc110l.pdf?ts=1591945883795&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FCC110L

Ralf9, 2020-06-12:
Dies fehlt dem CC110L gegenüber dem cc1101:
•Forward error correction (FEC) and interleaving
•Wake-on-Radio (WOR) and RCOSC (no RX timeout possible)
•Data Whitening
•Preamble quality threshold (PQT) indication (used to gate sync word detection)
•Link quality (LQI) indication (received signal strength (RSSI) is supported in Value Line parts)
•Temp sensor
•PA ramping and shaping
•MSK and ASK modulation (OOK modulation is supported in Value Line parts)
•No pin control for strobe commands (SPI strobe commands need to be used)
•ATEST removed

elektron-bbs, 2020-06-12:
Da müssen wir noch 0x17 für den CC110L einbauen. Im PDF SWRS109A war es noch 0x07.
@Ralf9 
Bei deiner Version 0x18 müsste es sich um einen CC113L handeln.

Ralf9, 2020-06-12:
Ja es ist der CC113L es gibt eine Version 0x08 und 0x18

fredlcore, 2020-06-12:
> Kannst Du mal versuchen den cc1xxx chip auf dem 433 Mhz Modul zu fotografieren oder abzulesen, falls er nicht vom Schrumpfschlauch verdeckt ist.
> Die Beschriftung auf dem 433 Mhz Modul ist auch interessant.

Klar:
[image]
[image]

Ich würde da draus lesen, dass es ein CC1101 ist - zumindest dem Aufdruck des Chips nach...

Ralf9, 2020-06-12:
ja der Beschriftung nach ist es ein CC1101.
Auf die Versionnr kann man sich nicht immer verlassen.
Ich habe laut Beschriftung und Versionsnr ein CC113L der laut Datenblatt keinen Sender hat, aber das Senden funktioniert trotzdem.

Du kannst mal ca 10 mal set raw C31 ausführen es muss immer das gleiche rauskommen.

fredlcore, 2020-06-12:
Done:
```
2020.06.13 00:28:16 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:16 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:16 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:16 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:17 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:17 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:19 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:19 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:19 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:19 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:19 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:19 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:21 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:21 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:21 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:21 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:22 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:22 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:23 4: SignalDuino: Read, msg READredu: MU;P0=-229;P1=249;P2=-486;P3=499;P4=-4880;P5=737;P6=-731;D=0121212301212303030301212123012123030121212123030121212121230301212121230123012121230121214565656561212123012121230121230303030121212301212303012121212303012121212123030121212123012301212123012121;CP=1;R=30;
2020.06.13 00:28:23 4: SignalDuino: Parse_MU, Fingerprint for MU Protocol id 27 -> EFTH-800 matches, trying to demodulate
2020.06.13 00:28:23 4: SignalDuino: Parse_MU, Decoded matched MU Protocol id 27 dmsg W27#113C4C306144 length 48 dispatch(1/4) RSSI = -59
2020.06.13 00:28:23 4: SignalDuino: SD_WS_Parse protocol 27, rawData 113C4C306144
2020.06.13 00:28:23 4: SignalDuino: SD_WS_27 Parse msg 113C4C306144, CRC 44
2020.06.13 00:28:23 4: SignalDuino: SD_WS_Parse decoded protocol-id 27 (EFTH-800), sensor-id 13C
2020.06.13 00:28:25 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:25 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:25 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:25 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:25 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:25 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:27 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:27 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:27 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:27 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:27 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:27 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:29 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:29 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:29 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:29 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:30 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:30 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:32 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:32 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:32 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:32 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:32 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:32 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:34 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:34 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:34 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:34 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:34 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:34 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:34 4: SignalDuino: Read, msg READredu: MU;P0=507;P1=-230;P2=258;P3=-485;P4=-4884;P5=744;P6=-718;D=0123232301232301230123232323230123230101230101012301232323232301010123230101232323012301010124565656562323012323230123230123012323232323012323010123010101230123232323230101012323010123232301230101012;CP=2;R=40;
2020.06.13 00:28:34 4: SignalDuino: Parse_MU, Fingerprint for MU Protocol id 27 -> EFTH-800 matches, trying to demodulate
2020.06.13 00:28:34 4: SignalDuino: Parse_MU, Decoded matched MU Protocol id 27 dmsg W27#22504DD07317 length 48 dispatch(1/4) RSSI = -54
2020.06.13 00:28:34 4: SignalDuino: SD_WS_Parse protocol 27, rawData 22504DD07317
2020.06.13 00:28:34 4: SignalDuino: SD_WS_27 Parse msg 22504DD07317, CRC 17
2020.06.13 00:28:34 4: SignalDuino: SD_WS_Parse decoded protocol-id 27 (EFTH-800), sensor-id 250
2020.06.13 00:28:36 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:36 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:36 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:36 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:36 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:36 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:37 4: SignalDuino: Set_raw, raw C31
2020.06.13 00:28:37 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:37 4: SignalDuino: SendFromQueue, called
2020.06.13 00:28:37 4: SignalDuino: Read, msg: C31 = 17
2020.06.13 00:28:38 4: SignalDuino: HandleWriteQueue, called
2020.06.13 00:28:38 4: SignalDuino: HandleWriteQueue, nothing to send, stopping timer
2020.06.13 00:28:41 4: SignalDuino: KeepAlive, ok, retry = 0
´``

fredlcore, 2020-06-12:
Aber ist denn die Tatsache, dass sich der Chip anders als erwartet meldet eine Erklärung für die Abstürze? Ich meine, entweder kann er das, was man von ihm verlangt, oder nicht, und das tut er ja ein paar Wochen auch...

sidey79, 2020-06-12:
Das mit dem unknown oder nicht, ist nur eine Anzeige.
Es hat mit der eigentlichen Funktion nichts zu tun.

Es ging ja eher darum, herauszufinden ob Du einen anderen chip hast.

fredlcore, 2020-06-12:
Ok.
Kann es denn ansonsten damit zu tun haben, dass die von mir verwendeten EFTH-800 Sensoren relativ neu implementiert wurden? Ich weiß nicht, auf welcher Ebene die Verarbeitung stattfindet, ob nun in der Nano-Firmware oder im FHEM-Modul, aber könnten malformed/unbekannte Pakete o.ä. ein Problem sein, das Abstürze a la buffer overflow auslöst?
Ich kenne so ein ähnliches Verhalten aus meinem Projekt, wo das Ethernet Shield nach ein paar Wochen abschmiert, aber Pings noch beantwortet werden. Der Arduino selber auf der seriellen Konsole aber weiterhin regulär ansprechbar ist. Der Grund da war dann, dass in zu kurzer Zeit zu viele Anfragen kamen und das Shield keine freien Sockets mehr hatte und dann auch nur noch das Stromlosmachen geholfen hat. Lässt sich sicher nicht 1:1 auf die Situation hier übertragen, aber irgendwie müsste sich das Problem ja einkreisen lassen...

Ralf9, 2020-06-12:
Dies sind komprimierte Mu Nachrichten, immer eine Nachricht mit 2 Wiederholungen.
Es sieht so aus, als würde dies so empfangen.
mit 
**set sduino raw CDR**
kannst Du die Komprimierung ausschalten. 
Wenn der Fehler wieder auftritt, bitte versuche ob Du mit
**set sduino raw XQ**
den Empfang abschalten kannst

```
02 M u ; b0 f0 80 ; 91 f7 81 ; 92 f1 80 ; b3 e0 81 ; a4 80 93 ; 95 e1 82 ; b6 d8 82 ; D 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 45 65 65 65 62 32 32 32 32 31 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 ; 43 32 ; 52 31 33 ; 03 0a 
02 M u ; b0 f3 80 ; 91 f7 80 ; b2 df 81 ; 93 ee 81 ; a4 90 93 ; 95 ea 82 ; b6 d4 82 ; D 01 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 23 01 23 03 03 01 23 01 23 01 45 65 65 65 61 21 23 01 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 23 01 23 03 03 01 23 01 23 01 ; 43 31 ; 52 32 42 ; 03 0a 
02 M u ; b0 fd 80 ; 91 f2 80 ; b2 e8 81 ; 93 e3 81 ; a4 88 93 ; 95 ea 82 ; b6 d7 82 ; D 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 45 65 65 65 61 21 21 23 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 ; 43 31 ; 52 31 42 ; 03 0a 

02 M u ; b0 ee 80 ; 91 e8 81 ; 92 f1 80 ; b3 e5 81 ; a4 a0 93 ; 95 da 82 ; b6 dd 82 ; D 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 45 65 65 65 62 32 32 32 32 31 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 ; 43 32 ; 52 31 32 ; 03 0a 
02 M u ; b0 ed 80 ; 81 81 81 ; b2 e6 81 ; 93 e8 81 ; b4 fc 92 ; 95 ea 82 ; b6 cf 82 ; D 01 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 23 01 23 03 03 01 23 01 23 01 45 65 65 65 61 21 23 01 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 23 01 23 03 03 01 23 01 23 01 ; 43 31 ; 52 32 41 ; 03 0a 
02 M u ; b0 e8 80 ; 91 f5 80 ; b2 e4 81 ; 93 ef 81 ; a4 9c 93 ; 95 e3 82 ; b6 d8 82 ; D 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 45 65 65 65 61 21 21 23 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 ; 43 31 ; 52 31 43 ; 03 0a 

02 M u ; b0 fb 80 ; 91 e2 81 ; 92 ee 80 ; b3 f1 81 ; a4 80 93 ; 95 e0 82 ; b6 d9 82 ; D 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 45 65 65 65 62 32 32 32 32 31 01 02 32 32 31 01 02 32 32 32 32 31 02 32 31 02 31 01 02 32 32 32 32 32 32 32 32 31 01 02 32 31 02 31 01 01 01 02 32 31 02 32 32 ; 43 32 ;52 31 33 ; 30
02 M u ; b0 e5 80 ; 91 ff 80 ; b2 e0 81 ; 93 f4 81 ; b4 fc 92 ; 95 ec 82 ; b6 d3 82 ; D 01 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 23 01 23 03 03 01 23 01 23 01 45 65 65 65 61 21 23 01 21 21 23 01 21 23 01 23 01 21 21 21 21 23 01 21 23 03 01 23 01 21 21 21 21 21 21 21 21 23 03 03 01 23 01 23 01 23 03 03 0 23 01 23 01 ; 43 31 ; 52 32 41 ; 03 0a 
02 M u ; b0 f0 80 ; 91 f3 80 ; b2 e4 81 ; 93 f1 81 ; a4 8c 93 ; 95 eb 82 ; b6 d2 82 ; D 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 45 65 65 65 61 21 21 23 01 21 21 23 01 21 23 03 03 03 01 21 21 23 01 21 23 01 23 03 03 03 01 23 01 21 21 21 21 23 03 01 21 21 21 21 23 03 03 01 21 21 21 21 21 ; 43 31 ; 52 31 42 ; 03 0a 
```

fredlcore, 2020-06-13:
Nur zum Verständnis:
Die Komprimierung soll ich jetzt schon abschalten und falls der Fehler wieder auftritt nach dem Ab- und wieder Anstecken (vorher kann ich dann ja keine Befehle mehr absetzen) den Empfang abschalten? 
Das würde dann ja bedeuten, dass ich dann den Adapter 3-4 Wochen ohne Datenempfang laufen lassen muss, um zu wissen, dass der Fehler daran liegt. Und wenn es daran nicht liegen sollte, dann ebenso wieder bei der nächsten Hypothese - da wären wir ja Monate mit der Fehlersuche beschäftigt...

Ralf9, 2020-06-13:
Die Bin Daten sind nach Ascii gewandelt komprmierte Mu Nachrichten, der Empfang funktioniert demnach ganz normal weiter und es können "nur" keine Befehle mehr an den nano gesendet werden.
Wenn der reset über USB nicht mehr funktioniert, kann sich auch der FTDI teilweise aufgehangen haben.
Hast Du schon getestet ob der USBreset im normalbetrieb funktioniert? Dies erkennst Du u.a. an der zurückgesetzten uptime.
Bei öffnen des serial Monitors von der Arduino IDE wird auch ein USBreset ausgeführt.

Als das Senden der Befehle nicht mehr funktioniert hat, hast Du da darauf geachtet ob noch bei jedem Befehl die LED geblinkt hat?

fredlcore, 2020-06-13:
Werde ich prüfen, wie frage ich die Uptime des Nano ab?

Ralf9, 2020-06-13:
> wie frage ich die Uptime des Nano ab

Mit dem raw Befehl **t**

Es kann auch sein, daß das senden eine Befehls schon früher nicht mehr funktioniert.
Wenn Du mit dem sduino keine Steckdosen schaltest, kann es eine Weile dauern bis es auffällt

Wenn 1 Minute nichts empfangen wird sendet die Keepalive Routine ein Ping, wird mehrere Minuten nichts empfangen, wird zuerst ein Reset versucht und ein "set close" ausgeführt

fredlcore, 2020-06-13:
Hm, mit 
```
get SignalDuino raw t
```
bekomme ich ein leeres Feld in FHEM zurück...

Ralf9, 2020-06-13:
Im fhem Modul gehts mit
**get uptime**
in einem seriellen Terminal wie z.B. dem Seriell Monitor der Arduino IDE ist der Befehl **t**

fredlcore, 2020-06-13:
Danke, seltsamerweise bringt mir get SignalDuino uptime nichts zurück, aber wenn ich in minicom ein Hangup auslöse, und dann "t" eingebe, sehe ich, dass die Uptime zurückgesetzt wird.
Soll ich dann zusätzlich noch den Reset-Button ausprobieren oder ist damit die Funktionsweise gesichert?

Ralf9, 2020-06-13:
es gibt im get Menü ein Auswahlfeld "uptime" 

Wenn sich der nano wieder aufgehangen hat, kannst Du auch mal den Reset-Button testen

fredlcore, 2020-06-13:
Ja, da kommt auch nichts, das ist ja im Endeffekt genau das gleiche, ob ich nun das Auswahlmenü bei "get" nehme oder oben in der Kommandozeile get SignalDuino uptime eingebe.

Ralf9, 2020-06-13:
bitte teste auch mal
**set sduino raw t**
das Ergebins müsste dann im log stehen

fredlcore, 2020-07-11:
Sorry für die längere Pause, ich wollte jetzt mal abwarten, wann er wieder crasht, das war jetzt am 4. Juli, so dass er dieses Mal fast vier Wochen (seit 13. Juni) durchlief.
Ein Wiederherstellen der Verbindung über set reset hat ebenso wenig gebracht, wie das reine Öffnen einer Terminalverbindung:
```
2020-07-05_12:13:38 no SIGNALduino found
2020-07-05_12:13:38 closed
2020-07-11_08:13:24 reset
2020-07-11_08:13:36 no SIGNALduino found
2020-07-11_08:13:36 closed
2020-07-11_08:15:53 reset
2020-07-11_08:16:05 no SIGNALduino found
2020-07-11_08:16:05 closed
2020-07-11_08:16:37 reset
2020-07-11_08:16:39 opened
```

Allerdings hat es diesmal funktioniert, ihn wieder anzusprechen, wenn ich im Terminal explizit einen Hangup ausgelöst habe (das sind die letzten beiden Zeilen oben).

Die Frage ist für mich jetzt, inwiefern sich ein Hangup über minicom von einem set reset unterscheidet und ob man das irgendwie abbilden könnte. Dann könnte ich in FHEM den state überwachen, und wenn der auf "closed" geht, entsprechend einen Reset auslösen. Gibt es dazu eine Möglichkeit?

sidey79, 2020-07-11:
Danke für die Information.
Offensichtlich gibt es hier Unterschiede die zu untersuchen wären.

Wie wir die Ursache für den Crash finden würde mich aber dennoch sehr interessieren.

fredlcore, 2020-07-11:
Im normalen Log ist leider nichts auffälliges zu finden. Wenn ich etwas tun kann, um etwas zur Klärung beizutragen, dann gerne. 
Gibt es denn sonst eine Möglichkeit herauszufinden, worin der Unterschied zwischen Minicom hangup und "set reset" besteht, um das nachbilden zu können?

sidey79, 2020-07-11:
Auf die Schnelle habe ich keine Spezifikation von hangup gefunden.

Beim Radino haben wir einen Fall, dass wir eine spezielle Reset Sequenz einhalten müssen um ihn in den Flash Modus zu versetzen. Das hatten wir in Perl leider auch nicht nachstellen können und über ein shellcommando realisiert.
Das wäre hier vielleicht auch eine Option, dass wir über die Shell einen Reset versuchen, wenn alles fehl schlägt.

Was den eigentlich Fehler angeht, könnte ein langzeitlog mit Verbose 5 unter Umständen helfen .
Eventuell lassen sich aus der letzten Seriellen Übertragung noch Rückschlüsse ziehen.
Ich hatte auch schon in der Vergangenheit spezielle Debug Firmwares gebaut.
Am Ende wird das alles dauern, weil ich ja aktuell noch nicht einmal einen Hauch einer Ahnung habe von welchem Programmteil der Fehler stammt.

fredlcore, 2020-07-11:
Ok, dann versuche ich es mal mit verbose 5 und hoffe, meine Logfiles quellen nicht über ;)...

HomeAutoUser, 2020-07-11:
@sidey79 
Ich komme hier an den Punkt erneut, ob man eine Funktion umsetzt, das sämtliche Ausgaben vom uC bei verbose 5 in ein anderes log gehen würden. Das würde dem User einiges erleichtern und auch hinterher. Wie denkst du darüber?

sidey79, 2020-07-11:
Das ist bereits über filelog möglich

HomeAutoUser, 2020-07-11:
Bitte nicht falsch verstehen.
Die uC Ausgaben umleiten in ein Filelog welches nicht in das Standard FHEM Logfile schreibt. Sozusagen parallel 2 Logs laufen lassen.

Wenn das schon geht, wie das?

sidey79, 2020-07-12:
Das Attribut nennt sich:

**eventlogging**
With this attribute you can control if every logmessage is also provided as event. This allows to generate event for every log messages. Set this to 0 and logmessages are only saved to the global fhem logfile if the loglevel is higher or equal to the verbose attribute. Set this to 1 and every logmessages is also dispatched as event. This allows you to log the events in a seperate logfile.

Es wird halt Zusätzlich noch weiter geloggt.

fredlcore, 2020-07-12:
So, das hat diesmal leider (bzw. zum Glück für das Logfile) nicht so lange gehalten, hier das FHEM-Logfile gefiltert nach den SignalDuino-Einträgen im Anhang bzw. die letzten Zeilen hier zum Mitlesen:
```
2020.07.11 18:17:30 5: SignalDuino: Parse_MU, start pattern for MU Protocol id 27 -> EFTH-800 not found, aborting
2020.07.11 18:17:30 5: SignalDuino: Read, RAW rmsg: Mu;∞˚Ä;ë–Å;íÒÄ;≥‰Å;§òì;ï€Ç;∂‹Ç;ßêı;D22122221211122222111211Eeeeb2222122122221211122222111211q;C2;R6;
2020.07.11 18:17:30 4: SignalDuino: Read, msg READredu: MU;P0=-251;P1=464;P2=241;P3=-484;P4=-4888;P5=731;P6=-732;P7=-29968;D=01023232310102323232323102323102310101023102323232323231023102310232310101023101010102456565656232323232310102323231010232323232310232310231010102310232323232323102310231023231010102310101010271;CP=2;R=6;
2020.07.11 18:17:30 4: SignalDuino: Parse_MU, Fingerprint for MU Protocol id 27 -> EFTH-800 matches, trying to demodulate
2020.07.11 18:17:30 5: SignalDuino: Parse_MU, part is 232323232310102323231010232323232310232310231010102310232323232323102310231023231010102310101010 starts at position 0 and ends at 104
2020.07.11 18:17:30 5: SignalDuino: Parse_MU, Starting demodulation (StartStr: 56565656 first found at 87 regex: (?:56565656)((?:10|23){48,}) Pos 87) length_min_max (48..48) length=48
2020.07.11 18:17:30 5: SignalDuino: Parse_MU, dispatching hex: W27#06304BA054EF
2020.07.11 18:17:30 4: SignalDuino: Parse_MU, Decoded matched MU Protocol id 27 dmsg W27#06304BA054EF length 48 dispatch(1/4) RSSI = -71
2020.07.11 18:17:30 5: SignalDuino: Dispatch, W27#06304BA054EF, test ungleich: disabled
2020.07.11 18:17:30 5: SignalDuino: Dispatch, W27#06304BA054EF, -71 dB, dispatch
2020.07.11 18:17:30 5: SignalDuino: dispatch W27#06304BA054EF
2020.07.11 18:17:30 4: SignalDuino: SD_WS_Parse protocol 27, rawData 06304BA054EF
2020.07.11 18:17:30 4: SignalDuino: SD_WS_27 Parse msg 06304BA054EF, CRC ef
2020.07.11 18:17:30 4: SignalDuino: SD_WS_Parse decoded protocol-id 27 (EFTH-800), sensor-id 630
2020.07.11 18:17:32 5: SignalDuino: Read, RAW rmsg: Mu;†òÇ;ëÿÜ;í∑É;£°Ö;§úó;d111211B122222

sidey79, 2020-07-12:
Ich werde mir diesen Teil mal ansehen.

`
2020.07.11 18:17:32 5: SignalDuino: Read, RAW rmsg: Mu; ˜‚;‘Ø†;’·ƒ;£¡…;¤œ—;d111211B122222 ;C2;RE0;
2020.07.11 18:17:32 4: SignalDuino: Read, msg READredu: MU;P0=-536;P1=1752;P2=951;P3=-1313;P4=-5916;D=0101023101010231010231023231023142310232323232320;CP=2;R=224;`
`

fredlcore, 2020-07-12:
Danke!

sidey79, 2020-07-13:
Leider habe ich da erst einmal keinen Fehler feststellen können. :(

An dem Reset Thema habe ich leider auch nicht viel gefunden wie die sequenz aussieht

fredlcore, 2020-07-13:
Hier ist zumindest der Hangup Code aus der functions.c aus dem Minicom Sourcecode:
```
/* Hangup. */
void m_hangup(void)
{
  int sec = 1;

  if (isdigit(P_MDROPDTR[0]))
    sscanf(P_MDROPDTR, "%2d", &sec);

  if (P_MDROPDTR[0] == 'Y' || (isdigit(P_MDROPDTR[0]) && sec > 0)) {
    m_dtrtoggle(portfd, sec);  /* jl 22.06.97 */
  } else {
    m_puts(P_MHANGUP);
    sleep(1);
  }
  /* If we don't have DCD support fake DCD dropped */
  bogus_dcd = 0;
}
```

fredlcore, 2020-07-14:
Hier noch der detaillierte Code, ggf. reicht das temporäre setzen der Übertragungsgeschwindigkeit auf 0 und dann wieder auf 57600 aus...
```
   79 /*
   80  * Drop DTR line and raise it again.
   81  */
   82 void m_dtrtoggle(int fd, int sec)
   83 {
   84 #ifdef USE_SOCKET
   85   if (portfd_is_socket)
   86     return;
   87 #endif
   88   {
   89 #ifdef TIOCSDTR
   90     /* Use the ioctls meant for this type of thing. */
   91     ioctl(fd, TIOCCDTR, 0);
   92     if (sec>0) {
   93       sleep(sec);
   94       ioctl(fd, TIOCSDTR, 0);
   95     }
   96 
   97 #else /* TIOCSDTR */
   98 #  if defined (POSIX_TERMIOS)
   99 
  100     /* Posix - set baudrate to 0 and back */
  101     struct termios tty, old;
  102 
  103     tcgetattr(fd, &tty);
  104     tcgetattr(fd, &old);
  105     cfsetospeed(&tty, B0);
  106     cfsetispeed(&tty, B0);
  107     tcsetattr(fd, TCSANOW, &tty);
  108     if (sec > 0) {
  109       sleep(sec);
  110       tcsetattr(fd, TCSANOW, &old);
  111     }
  112 
  113 #  else /* POSIX */
  114 #    ifdef _V7
  115 
  116     /* Just drop speed to 0 and back to normal again */
  117     struct sgttyb sg, ng;
  118 
  119     ioctl(fd, TIOCGETP, &sg);
  120     ioctl(fd, TIOCGETP, &ng);
  121 
  122     ng.sg_ispeed = ng.sg_ospeed = 0;
  123     ioctl(fd, TIOCSETP, &ng);
  124     if (sec > 0) {
  125       sleep(sec);
  126       ioctl(fd, TIOCSETP, &sg);
  127     }
  128 
  129 #    endif /* _V7 */
  130 #  endif /* POSIX */
  131 #endif /* TIOCSDTR */
  132   }
  133 }
```

fredlcore, 2020-07-14:
In der 00_SIGNALduino.pm sehe ich, dass es da für $hardware eq "radinoCC1101" schon einen Fall gibt, wo dann über stty der Port geöffnet und geschlossen wird. Vielleicht wäre das für die Variante, die ich hier habe, auch der richtige Weg?

sidey79, 2020-07-14:
Daran habe ich auch schon gedacht.

leapfog, 2020-11-04:
Ich habe seit kurzer Zeit (vermutlich) dasselbe Problem, allerdings tritt es früher auf: Nach etwa vier Stunden reagiert plötzlich mein SIGNALduino nicht mehr. Weder Senden noch Empfangen. Der Status in FHEM steht auf **closed**. Bisher kann ich das Problem nur dadurch beheben, dass ich den USB-Stick heraus ziehe, kurz warte und ihn dann wieder einstecke.

Da es weiter oben abgefragt wurde: **lsusb** zeigt ihn als **1da9:002c Unknown radino CC1101**, ein **set sduino raw C31** bringt **Read, msg: C31 = 14**. Wenn ich mich recht erinnere leuchtet die grüne LED nahe dem USB-Anschluss dann weiterhin grün, die anderen LEDs sind aus.

Der SIGNALduino hat >10 Monate hinter sich, in denen er tadellos funktioniert hat. Ich betreibe ihn nicht direkt am FHEM Rechner sondern an einem zweiten via **socat/ser2net**. Kürzlich ist mir aufgefallen, dass auf dem Stick nicht die neuste Firmware lief. Ich dachte ich hätte gelesen, dass FHEM den SIGNALduino automatisch aktualisiert und vermutete das geht wegen dem socat/ser2net nicht. Also habe ich den Stick an meinen Desktop angeschlossen, den Reset-Taster zweimal gedrückt und mit avrdude die 3.4.0 geflasht.

Kurz danach traten die Probleme zum ersten Mal auf. Gestern bin ich zurück auf die 3.3.1 gegangen, das hat aber mein Problem offenbar nicht behoben. Wie weiter oben angeregt habe ich gerade eben mit **set sduino raw CDR** die Komprimierung abgeschaltet. Bleibt das so, auch wenn ich den Stick stromlos mache? Oder muss ich es danach erneut ausschalten?

HomeAutoUser, 2020-11-04:
Hallo @leapfog 
das Verhalten beim Radino kann ich leider nicht bestätigen.
Ich betreibe ihn seit längerem problemlos mit der Firmware welche bei dir abstürzt.

Hast du mal `raw e` probiert dor das Verhalten wenn du `raw CDR` ausgeführt hast?

leapfog, 2020-11-04:
Die Firmware mit der ich ihn gekauft habe (vermutlich 3.3.1) hat >10 Monate ohne Abstürze funktioniert. Aber da es immer wieder Meldungen über _unbekannte Protokolle_ gab hatte ich die Hoffnung ein Wechsel auf 3.4.0 würde das verbessern. Nachdem er dann regelmäßig innerhalb weniger Stunden abgestürzt ist bin ich zurück auf die 3.3.1. Dummerweise sind die Abstürze geblieben.

Als ich den Stick am Wochenende zum Flashen an meinem Desktop hatte bin ich mal mit **tio** dran gegangen und habe **V**, **t** und **e** ausprobiert. Ich weiss aber nicht mehr, ob das war bevor oder nachdem ich wieder auf die ältere Firmware zurück gegangen bin.

Nach dem **raw CDR** heute Vormittag hat der Stick ca. 10 Stunden durchgehalten. Danach nur noch etwa eine Stunde. Ich vermute ich hätte die Kompression erneut ausschalten müssen. Oder das hatte gar keinen Einfluss. Ich ziehe ihn jetzt wieder raus, stecke ihn wieder rein, mache ein **raw e** und ein **raw CDR**. Mal sehen wie lange er dann durchhält.

Ach ja, ich habe nachgeprüft: Wenn der Stick nicht mehr reagiert ist (wie oben beschrieben) die grüne LED nahe dem USB-Anschluss dauerhaft grün und die anderen LEDs sind dauerhaft aus.

sidey79, 2020-11-04:
Der CDR Befehl wird im eeprom gespeichert.
Nur, wieso lässt Du nicht alles im Standard?

leapfog, 2020-11-04:
Ich habe den Stick >10 Monate lange so genutzt wie ich ihn gekauft habe. Ohne etwas zu verändern. Alles lief zufriedenstellend, bis auf gelegentlich auftretenden Meldungen, dass manche Signale nicht erkannt wurden.

Da hin und wieder die Meldung kam, ich solle dies an den Entwickler melden, dachte ich, vorher sollte ich besser mal die neuste Version der Firmware installieren. Vielleicht erkennt die ja diese Signale inzwischen.

Nach dem Upgrade auf die 3.4.0 ist der Stick dann irgendwann abgestürzt (und ich habe es erst nach ~24 Stunden gemerkt). Als erstes habe ich alle beteiligten Komponennten einmal neu gestartet habe und auch ein paar der täglichen Updates von FHEM selbst abgewartet.

Aber der Stick ist weiterhin alle paar Stunden abgestürzt. Deswegen habe ich die 3.3.1 wieder drauf geflashet, aber das hat das Problem auch nicht gelöst.

Also habe ich die issues überflogen und glaubte dieses hier würde mein Problem beschreiben. Also habe ich die hier vorgebrachten Fragen und Anweisungen `Ralf9 am 12.Juli: bitte versuch mal ein "set sduino raw C31"` und `Ralf9 am 13 Jun: mit "set sduino raw CDR" kannst Du die Komprimierung ausschalten.` ausgeführt und dokumentiert.

HomeAutoUser, 2020-11-05:
Hallo @leapfog ,
ist das Absturzverhalten reproduzierbar wennn du jetzt deinen Empfänger vom Netz trennst und dann wieder ansteckst?
Wenn ja, so würde ich mal vorschlagen, das du ihn ansteckst und vom FHEM "abkoppelst" (disconnectest) und nur via einem Terminalprogramm auf ihn schaust. Wenn der Absturz erfolgen sollte, so sollte im Terminal die letzt empfangene Nachricht zu sehen sein. Diese mal speichern und den Vorgang wiederholen. Diese Nachrichten dann hier posten. Vielleicht ist ein "spezielles" Signal bei dir im Empfang, was die Hardware zum abstürzen bringt. Wie schon erwähnt, hier läuft die Hardware ohne Probleme durch.

LG

leapfog, 2020-11-06:
Wenn der SIGNALduino nicht mehr reagiert hat musste ich ihn immer vom Rechner trennen und wieder neu anstecken. Es ist ein USB-Stick, den habe ich immer rausgezogen, kurz gewartet und wieder eingesteckt.

Anfangs hatte ich die Abstürze nicht erwartet und deswegen nicht genau beobachtet. Als ich anfing die Zeit zu messen waren es mehrfach ca. vier Stunden Laufzeit bis zum Absturz. Inzwischen sind es mal zwei, mal zehn und einmal mehr als 15 Stunden. Länger als einen Tag hat er aber meines Wissens in den letzten zehn Tagen nicht durchgehalten.

Ich werde den Stick jetzt wieder mit der 3.4.0 flashen und dann (dauerhaft) mit einem Terminalprogramm laufen lassen.

HomeAutoUser, 2020-11-06:
> Als ich anfing die Zeit zu messen waren es mehrfach ca. vier Stunden Laufzeit bis zum Absturz. Inzwischen sind es mal zwei, mal zehn und einmal mehr als 15 Stunden. Länger als einen Tag hat er aber meines Wissens in den letzten zehn Tagen nicht durchgehalten.

Solch ein Verhalten könnte mit hoher Wahrscheinlichkeit ein Hardware / Firmware Problem sein. Wir sind mal auf die Terminalaufzeichnungen gespannt.

> Alles lief zufriedenstellend, bis auf gelegentlich auftretenden Meldungen, dass manche Signale nicht erkannt wurden.

Wie äußerte sich das bzw. was meinst du damit? Eine FHEM Meldung?

leapfog, 2020-11-06:
Hier ein paar Zeilen aus meinen /opt/fhem/log/fhem-2020-??.log Dateien:

2020.11.02 19:45:58 4: sduino: CUL_TCM97001 Unknown 61 (3D4105A1028000) length: 14
2020.11.02 19:50:57 4: sduino: CUL_TCM97001 Unknown 61 (3D4105A1028000) length: 14
2020.11.02 19:55:56 4: sduino: CUL_TCM97001 Unknown 61 (3D4105A10A0000) length: 14

2020.09.21 18:40:06 1: SD_UT Please report maintainer. Your model from Atlantic Security are unknown! rawData=567E4E0FF
2020.10.27 20:31:17 1: SD_UT Please report maintainer. Your model from Atlantic Security are unknown! rawData=000000000
2020.10.28 22:25:41 1: SD_UT Please report maintainer. Your model from Atlantic Security are unknown! rawData=0B6F230A9

2020.07.23 23:38:08 1: sduino: SD_UT_Parse UNDEFINED sensor unknown detected, protocol 34, data 00000, code 000
2020.10.30 18:19:29 1: sduino: SD_UT_Parse UNDEFINED sensor unknown detected, protocol 34, data 4FFAF, code 4FF
2020.10.30 18:23:02 1: sduino: SD_UT_Parse UNDEFINED sensor unknown detected, protocol 34, data 4FFAF, code 4FF
2020.10.30 18:23:14 1: sduino: SD_UT_Parse UNDEFINED sensor unknown detected, protocol 34, data 9FF5E, code 9FF

2020.03.02 21:12:56 3: sduino: Unknown code 241B905AF140, help me!
2020.08.10 17:57:36 3: sduino: Unknown code 246F8157F210, help me!
2020.10.20 14:29:00 3: sduino: Unknown code 247890FFF120, help me!

2020.05.17 10:13:41 3: sduino: Unknown code i19915AF51100, help me!
2020.10.02 01:39:05 3: sduino: Unknown code i197059C80100, help me!
2020.10.28 09:15:58 3: sduino: Unknown code i195058F82100, help me!

2020.07.15 08:43:33 3: sduino: Unknown code P88#794FC60F9291A8424, help me!
2020.08.29 18:19:43 3: sduino: Unknown code P88#182797E29291A8424, help me!
2020.09.13 02:54:25 3: sduino: Unknown code P88#D3371AA69291A8424, help me!
2020.10.12 20:38:10 3: sduino: Unknown code P88#732F80579291A8424, help me!

2020.09.03 14:52:31 3: sduino: Unknown code u22#BBC77F20, help me!
2020.09.07 11:54:56 3: sduino: Unknown code u22#BBC27F53, help me!
2020.09.09 03:20:40 3: sduino: Unknown code u22#BBC57F70, help me!

2020.05.23 06:18:21 3: sduino: Unknown code u31#01B119, help me!
2020.06.18 09:16:08 3: sduino: Unknown code u31#01B71F, help me!
2020.07.04 07:57:23 3: sduino: Unknown code u31#01C028, help me!

2020.08.16 20:43:03 3: sduino: Unknown code u40#4360, help me!
2020.08.20 00:48:09 3: sduino: Unknown code u40#44300D54A, help me!
2020.08.20 04:06:33 3: sduino: Unknown code u40#446, help me!

2020.01.31 17:06:02 3: sduino: Unknown code u52#FFEA2F411E8253C87FF517A08F4129E43FFA8BD0002A2F411E8253C87FF517A08F4129E43FFA8BD00F4129E43, help me!
2020.01.31 17:42:25 3: sduino: Unknown code u52#FFEA2F411E4257B87FF517A08F212BDC3FFA8BD0002A2F411E4257B87FF517A08F212BDC3FFA8BD0479095EE1, help me!
2020.01.31 17:44:23 3: sduino: Unknown code u52#FFEA2F411E4257B87FF517A08F212BDC3FFA8BD0002A2F411E4257B87FF517A08F212BDC3FFA8BD0479095EE1, help me!

Ich hatte die Hoffnung durch das Firmware-Upgrade würden das weniger.

leapfog, 2020-11-07:
erster Crash, die letzten fünf Meldungen (Timestamps vom Terminalprogramm):

[08:15:55] MS;P0=487;P1=-978;P2=-1932;P3=-3912;D=03010202020201010102010102010101010202020102020102020202020101010201010102;CP=0;SP=3;R=51;m0;

[08:15:55] MU;P0=13408;P1=-711;P2=751;P3=273;P4=-461;P5=500;P6=-235;P7=-960;D=012121213434345656563456343434563456563434563456565634563434563456563456563456565656345637212121213434345656563456343434563456563434563456565634563434563456563456563456565656345637212121213434345656563456343434563456563434563456565634563434
56345656345656;CP=3;R=239;O;

[08:15:55] MU;P0=273;P1=-458;P2=519;P3=-214;P4=-968;P5=762;P6=-705;D=01232323230123045656565601010123232301230101012301;CP=0;R=243;

[08:15:55] MU;P0=-461;P1=510;P2=-214;P3=272;P4=-944;P5=755;P6=-711;D=012301212301212301212121230123456565656303030121210121230301230121212301230301230121230121230121212123012301212121230123456565656303030121210121230301230121212301230301230121230121230121212123012345656565630303012121230123030301230121230301230121212
30123;CP=3;R=242;O;

[08:15:56] MU;P0=-456;P1=273;P2=514;P3=-215;P4=-945;P5=769;P6=-698;D=010231023231023231023232323102314565656561010102320231456565656101010232323102310101310231010231023231023231023232323102314565656561010102320231456565656101010232323102310101023102323101023102323231023101023102323102323102323232310231456565656101010
23232;CP=1;R=243;O;

sidey79, 2020-11-07:
Den Fehler zu finden ist in etwa wie eine Nadel im Heuhaufen zu suchen.
Ich betreibe selbst einen Arduino über ser2net und der läuft seit Jahren stabil.

Ich kann nicht ausschließen, dass irgendwo ein Fehler im Programm ist, der dieses Verhalten verursacht.

Allerdings hatte ich auch schon diverse Erfahrungen mit dem ein oder anderen ähnlichen Verhalten.
Insbesondere die Tatsache, dass Du den Fehler erst hast, seit dem Du eine neuere Firmware eingespielt hast lässt erahnen, dass das Problem nicht direkt im Code steckt, sonst würde ein downgrade den Fehler eleminieren.

Folgendes würde ich raten:
Den Arduino Flash und ggf. den eeprom komplett löschen. Sozusagen alles mit 0 überschreiben.

Der flash kann mit avrdude gelöscht werden. Um den eeprom komplett zu löschen braucht es ein Programm, welches das erledigt. Das musst Du vermutlich selbst compilieren für den Radio.

Ich würde annehmen, dass der Fehler nicht aus dem eeprom kommt und stattdessen nur durch den Flash verursacht wird.
Also Flash vielleicht drei mal löschen und das Gerät für ein paar Minuten Stromlos machen.
Anschließend kannst Du die Firmware neu einspielen. Gerne auch erst mal die alte.

leapfog, 2020-11-07:
zweiter Crash:

MS;P1=-3894;P2=493;P3=-979;P4=-1961;D=21232424242423232324232324232323232424242423242323242424242323232423232324;CP=2;SP=1;R=59;O;

MS;P0=493;P1=-976;P2=-1952;P3=-3896;D=03010202020201010102010102010101010202020201020101020202020101010201010102;CP=0;SP=3;R=59;O;m2;

MS;P0=491;P1=-970;P2=-1956;P3=-3896;D=03010202020201010102010102010101010202020201020101020202020101010201010102;CP=0;SP=3;R=59;m1;

MS;P0=491;P1=-970;P2=-1956;P3=-3896;D=03010202020201010102010102010101010202020201020101020202020101010201010102;CP=0;SP=3;R=59;m0;

MU;P0=-14352;P1=531;P2=-218;P3=273;P4=-444;P5=-938;P6=772;P7=-690;D=01234123434121212121212341234341234121234341234341234343567676767341212123412341234341234341234123434121212121212341234341234121234341234341234343567676767341212123412341234341234341234123434121212121212341234341234121234341234341234343567676767341212123;CP=3;R=241;O;

[29 more replies]