RFD-FHEM/SIGNALDuino #139: Device hang off after set raw W

HomeAutoUser, 2020-07-05:
Entweder Softwarefehler in der Umsetzung oder FW Fehler ...

## Expected Behavior
write Register value

## Actual Behavior
- Zustand das das Device hängen bleibt und nicht mehr reagiert.
-> Device muss abgesteckt und neu angesteckt werden.
--> RESET via FHEM geht ins "leere"

## Steps to Reproduce the Problem

Auf die Aussage von @sidey79 
`Man kann mehrere CW Befehle auf einmal senden.`
habe ich mal getestet.

  1. aus FHEM heraus via Setauswahl `set raw W0211,0322`
  2.
  3.

GEHT NICHT

Selbst via Terminal gelang es mir nicht.
Bitte prüfen und dieser Zusand sollte abgefangen werden.

## Specifications

  - Microcontroller: radio & nano 
  - Version (Firmware): 3.4.0-dev & 3.5.0-dev
  - Versionmodul (FHEM Module):

HomeAutoUser, 2020-07-06:
Ich probierte es heute nochmal durch.
1) Vorsicht, es muss geklärt werden ob es wirklich ein BUG ist oder nur eine ungünstige Konstellation der Registerwerte.
2) Das Einizige was ein Bug sein könnte, das nach dem anwenden des raw Befehles e die Register zurückgesetzt werden aber nicht der empfang losläuft. Erst nach einem Reset geschiet dies.

>Meiner Ansicht nach, wird auch ein einzelnes serielles Kommando nicht verarbeitet, da keine Antwort zurück kommt.

Welche Erkenntnis hattest du erhalten, was dich dazu führt, zu urteilen es ist ein Fehler.

sidey79, 2020-07-06:
> > Meiner Ansicht nach, wird auch ein einzelnes serielles Kommando nicht verarbeitet, da keine Antwort zurück kommt.
> 
> Welche Erkenntnis hattest du erhalten, was dich dazu führt, zu urteilen es ist ein Fehler.

` ``2020.07.05 23:01:51.734 5: sduino: Parse, noMsg: W0211
2020.07.05 23:01:51.725 4: sduino: Read, msg: W0211
2020.07.05 23:01:51.705 5: sduino: SimpleWrite, W0211#W0322```

Bei dem aktuellen commit fehlt die Rückmeldung des Kommandos, welches verarbeitet wurde.
Das lässt schließen, dass die Verarbeitung nicht mehr an diesen Punkt gelangt.

elektron-bbs, 2020-07-06:
Konkret getestet haben wir heute folgenden Befehl:

`set sduino raw W0202#W0303`

Das CC1101-Register 0x00 hatte anschließend den Wert 0x02 und Register 0x01 den Wert 0x03. Das Setzen mehrerer Register über den raw-Befehl scheint also zu funktionieren.
Welche Firmware-Version da nun gerade aktiv war, kann ich jetzt nicht mehr sagen.

sidey79, 2020-07-06:
Ich hab jetzt auch etliche commits durchprobiert.

Kein Fehler mehr da. Sehr komisch. Für heute ist Feierabend

sidey79, 2020-07-10:
Habe es mit dem heute aktuellsten commit einer compilierten Version auf dem nano probiert.
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/dev-3.4.0%2B200711

 Set_raw, raw W0202
und auch
 Set_raw, raw W0202#W0302
funktionieren

```
2020.07.11 01:03:41.542 4: sduino: HandleWriteQueue, nothing to send, stopping timer
2020.07.11 01:03:41.535 4: sduino: HandleWriteQueue, called
2020.07.11 01:03:41.252 5: sduino: Parse, noMsg: W0302
2020.07.11 01:03:41.245 4: sduino: Read, msg: W0302
2020.07.11 01:03:41.238 5: sduino: Parse, noMsg: W0202
2020.07.11 01:03:41.231 4: sduino: Read, msg: W0202
2020.07.11 01:03:41.210 5: sduino: SimpleWrite, W0202#W0302
2020.07.11 01:03:41.198 4: sduino: SendFromQueue, called
2020.07.11 01:03:41.188 4: sduino: HandleWriteQueue, called
2020.07.11 01:03:41.075 5: sduino: AddSendQueue, sduino: W0202#W0302 (1)
2020.07.11 01:03:41.074 4: sduino: Set_raw, raw W0202#W0302
2020.07.11 01:03:29.027 4: sduino: HandleWriteQueue, nothing to send, stopping timer
2020.07.11 01:03:29.017 4: sduino: HandleWriteQueue, called
2020.07.11 01:03:28.715 5: sduino: Parse, noMsg: W0202
2020.07.11 01:03:28.708 4: sduino: Read, msg: W0202
2020.07.11 01:03:28.689 5: sduino: SimpleWrite, W0202
2020.07.11 01:03:28.681 4: sduino: SendFromQueue, called
2020.07.11 01:03:28.673 4: sduino: HandleWriteQueue, called
2020.07.11 01:03:28.560 5: sduino: AddSendQueue, sduino: W0202 (1)
2020.07.11 01:03:28.559 4: sduino: Set_raw, raw W0202
```

HomeAutoUser, 2020-07-11:
Wenn das nun funktioniert, dann ist es gut :)

Das Einzige was ich bemängel, das diese Art Der Schreibweise (mit mehreren W Kommandos und  #)  nicht in der Commandref steht. Das wusstest nur du ;)

sidey79, 2020-07-11:
Ich würde sogar soweit gehen, dass die komplette Spezifikation der Kommandos nirgendwo steht :(

HomeAutoUser, 2020-07-11:
Woran liegt das wohl ;)

HomeAutoUser, 2020-07-15:
Ich würde den Fall als geschlossen betrachten.

>Ich würde sogar soweit gehen, dass die komplette Spezifikation der Kommandos nirgendwo steht :(

Wurden hier https://github.com/RFD-FHEM/RFFHEM/pull/869 mit ergänzt in der Commandref.

closed 2020-07-15 by HomeAutoUser