RFD-FHEM/SIGNALDuino #153: Vorgehen dokumentieren, wie ein Release erzeugt wird:

sidey79, 2020-12-14:
- [ ] Release erzeugen
- [ ] Namensgebung der Releases 
- [ ] Release notes

sidey79, 2020-12-14:
### 1.Neue Release erstellen
[image]

### 2. Name des Tags, Branch und Release Bezeichnung angeben:

[image]

### 3. Entwurf öffnen 
[image]

### 4.Release fertigstellen. Für nicht finale Versionen mit Datumsangabe prerelease auswählen und speichern:

[image]

### 5. Anschließend werden die Firmware Dateien compiliert und in dem Release abgelegt. Das dauert ca. 1 Minute.
Der Status kann unter Actions nachgesehen werden

[image]

sidey79, 2020-12-15:
@HomeAutoUser 

Kannst Du dir die Anleitung mal ansehen ob Du das so verstehst?
Ich pack dass dann noch an eine geeignete Stelle, wenn es verständlich ist. So kannst Du z.B. auch testen ob die Anpassung für den 8mhz micro für den leonardo passt:

https://github.com/RFD-FHEM/SIGNALDuino/tree/compile-micro8Mhz

HomeAutoUser, 2020-12-15:
@sidey79 
>Kannst Du dir die Anleitung mal ansehen ob Du das so verstehst?

ich denke deine Beschreibung ist verständlich und auch die Fenster / Button kommen mir bekannt vor.

>Ich pack dass dann noch an eine geeignete Stelle, wenn es verständlich ist. So kannst Du z.B. auch testen ob die Anpassung für den 8mhz micro für den leonardo passt:
>
>https://github.com/RFD-FHEM/SIGNALDuino/tree/compile-micro8Mhz

Gibt es aktuell schon eine Datei welche nach deiner aktuellen Anpassung hinterlegt ist oder müsste ich das live durchspielen mit https://github.com/RFD-FHEM/SIGNALDuino/tree/compile-micro8Mhz und ein Pre-Release erstellen.

sidey79, 2020-12-15:
Ich habe kein Release auf Basis https://github.com/RFD-FHEM/SIGNALDuino/tree/compile-micro8Mhz erstellt.

Probier das gerne aus, Du kannst es dann ja auch wieder löschen.

HomeAutoUser, 2020-12-15:
Ich denke zurech zu kommen mit der Automatik. 
Soeben habe ich mal zum testen https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.5.1 generiert um die Files dann mit der Hardware zu testen.

Fragen:
1) 
Den Zwischenstand als du das eingebunden hast, können wir doch als Releasepoint bestimmt löschen, oder?
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.5.0
2) Kann man die Anpassungen dann auch soweit treiben, das im FHEM WEB unter set flash, nach einem get availableFirmware der neue Punkt erscheint? Im Testfall fehlt ja der Punkt `3.5.1-dev+20201215`

sidey79, 2020-12-15:
Ja das Tag können wir wieder entfernen

HomeAutoUser, 2020-12-15:
Achtung, die automatische Compilierung beinhaltet noch einen Fehler, welcher sich auch auf andere Hardware (bei mir ein Nano) ausweitet. Das hat mit dem Radino nichts zu tun.

Ich versuche dies noch weiter einzugrenzen.

sidey79, 2020-12-15:
Schade, dass es nicht klappt.
An dem cc1101 flag kann es dann ja wohl nicht liegen, wenn wir einen Dauersender gebaut haben

HomeAutoUser, 2020-12-15:
Ich kann mir nicht vorstellen, das es online nicht geht aber lokal schon. Leider weiß ich nicht den vollständigen Hintergrund aller online ablaufenden Aktionen.

sidey79, 2020-12-15:
Ist im Endeffekt nur ein einziger Befehl mit dem compilierte wird

HomeAutoUser, 2020-12-15:
Scheinbar muss sich dort ein Fehler einschleichen. Ich weiß nicht wie man dies weiter testen könnte.

sidey79, 2020-12-15:
@HomeAutoUser 

Wie genau hast Du festgestellt dass dauerhaft gesendet wird?
Ich habe die Firmware für nano328 mit und ohne cc1101 compiliert.

Ich konnte hier zunächst keinen Fehler feststellen.

HomeAutoUser, 2020-12-16:
@sidey79 
Ich / wir haben das Verhalten mitbekommen als ich nach dem fertigstellen des ReleasePoints die Firmware via aviableFirmware / dann flash erneuerte.

Als das Auftreten von keinem Empfang auftrat sind wir näher aktiv geworden. Fehler eingegrenzt das es den Dauersender hervorruft.

Firmware wieder downgegraded ging. Somit die Firmware im Verdacht gehabt. Alles probiert

1) Firmware selbst compiliert mit Sourcen (WIndows) —> lief Feherfrei

2) Firmware via get aviableFirmware / flash geladen —> Dauersender

3) Firmware als hex-File vom Releasepoint Assent Sammlung selbst herunter geladen und via lokalem Flash Befehl  auf Hardware gebracht —> Dauersender

4) offener Versuch, Sourcen mit Linux und Arduino compilieren

Ich kann es mir nur so erklären, das der verwendete Compiler der Action etwas missversteht.

sidey79, 2020-12-16:
Betrifft das nur die einfach Variante oder auch die mit cc1101?

Ich habe beide genauso compiliert, wie es die actions machen.
Compilier etc. habe ich identisch gewählt.
Konnte zunächst nichts feststellen, wo es denn liegen könnte.

HomeAutoUser, 2020-12-16:
Wir haben uns erstmal auf den Nano nur bezogen weil er einen weitere Nanos stört + ESP.

Wichtig ist, selbst compilierte Files (also selbst lokal comilieren lassen) funktionieren.
Die Hex-Files von hier 
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.5.0-dev%2B20201207
aus dem Assets eingespielt gehen nicht.

Hast du den Flash Vorgang mal so absolviert.

Bei einem Nano mit cc1101 habe ich es vergessen gestern Abend zu testen wie das Verhalten da ist und leider habe ich keinen cc1101 hier zur Hand.

sidey79, 2020-12-16:
Ich habe noch schnell einen gelötet, ich teste jetzt mal mit nano simple und nano cc1101 einen esp mit cc1101 habe ich hier auch noch liegen. Mal schauen, wie weit ich komme.
Irgendwo fliegt auch ein promicro rum

sidey79, 2020-12-16:
@HomeAutoUser 

Ich habe auf einem Nano mit einer LED an Pin D11 getestet:

Sowohl die Firmware
https://github.com/RFD-FHEM/SIGNALDuino/releases/download/3.5.0-dev%2B20201207/SIGNALDuino_nano328simple_3.5.0-dev+20201207.hex

Also auch diese:
https://github.com/RFD-FHEM/SIGNALDuino/releases/download/test2/SIGNALDuino_nano328simple_test2.hex

Und eine lokal compilierte aus dem Branch compile-micro8mhz.

Ich kann an dem Verhalten des Senders, also an der LED keinerlei Fehler feststellen.
Wenn ich `SR;R=20;P0=-30000;P1=30000;D=1010;` eingebe habe ich immer das gleiche Verhalten, dass die LED blinkt wenn gesendet ist. Vorher und nachher ist sie ausgeschaltet.
Wie hattest Du / Ihr getestet?

HomeAutoUser, 2020-12-16:
@sidey79 
den Test den ich machte mit einem Nano + LOW Budget Empfänger erfolgte so.

- get availableFirmware
- set flash 3.5.0-dev+20201207
- danach war der Dauersender vorhanden (dies haben wir auch am PIN gemessen) @elektron-bbs war da selbt tätig ;-)

Das Verhalten hatte ich mindestens 2x nachgestellt.

Hier habe ich soeben den Nano mit cc1101 geflasht und habe bisher keine Einschränkungen.

Das Verhalten mit dem Nano + LOW Budget kann ich frühestens morgen wieder testen.
Sehr sehr komisch auf jedenfall der Sachverhalt.

Kannst du den Re-leasePoint 3.5.0-dev+20201207 wieder "aktivieren" das wir diesen bei set flash sehen?

PS: das File `SIGNALDuino_leonardocc1101_3.5.0-dev+20201207.hex` geht nicht mit dem radino

sidey79, 2020-12-16:
@HomeAutoUser 

Okay das habe ich nachgestellt. 
Hardware auf `nano328 `gesetzt und dann über das Menü geflasht.
Er flasht die flasche Firmware und tatsächlich ist PIN#11 jetzt HIGH und die LED leuchtet.

`2020.12.16 18:24:12.313 3: sduino: Set_flash, filename FHEM/firmware/SIGNALDuino_nano328cc1101_3.5.0-dev+20201207.hex provided, trying to flash`

Das liegt an den veränderten Dateinamen :(

Das mit dem Leonardo brauchst Du nicht testen. Interessant ist, ob der micocc1101 von https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/test2 geht :)

elektron-bbs, 2020-12-16:
Ich habe an meinem SIGNALduino Nano ohne CC1101 schon immer eine LED an Pin 11 zur Kontrolle des Sendens.
Ich habe vor 2 Minuten die gleichen Befehle wie @HomeAutoUser ausgeführt. Die Firmware-Version ist jetzt:
V 3.5.0-dev+20201207 SIGNALduino  - compiled at Dec 14 2020 22:50:44

Die Sende-LED leuchtet dauernd!

HomeAutoUser, 2020-12-16:
@sidey79 
>Er flasht die flasche Firmware und tatsächlich ist PIN#11 jetzt HIGH und die LED leuchtet.

Schön, das wir nun den Fall nachstellen konnten und ebenso @elektron-bbs diesen Fall bestätigt.

>Das liegt an den veränderten Dateinamen :(

Dann muss da eine Anpassung her ;-)

>Das mit dem Leonardo brauchst Du nicht testen. Interessant ist, ob der micocc1101 von https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/test2 geht :)

Ich habe es soeben ausprobiert und leider NEGATIV, dies geht nicht. (Testfile: SIGNALDuino_microcc1101_test2.hex)
Da müssen wir bzw. sollten wir versuchen die orginal Sourcen einzubinden. Laut Doku sollte das auch gehen und wir haben ja auch ebenso die Sourcen für den ESP8266 oder ESP32 ... angegeben.

sidey79, 2020-12-16:
Okay, habe den Dateinamen eben fix angepasst.
Ist schon ein bisschen blöde die passende Version für den uC zu finden. Das geht nur über einen Teil des Dateinamens.

HomeAutoUser, 2020-12-16:
>Ist schon ein bisschen blöde die passende Version für den uC zu finden. Das geht nur über einen Teil des Dateinamens.

Das sollte uns damit sagen, das wir auf jedenfall den Dateinamen passend zum uC wählen und nach einem klarem Schema verfahren sollten bzw. Trennzeichen beinhalten sollten.

Wenn das Problem dann gefixt wurde, dann wäre das mit dem Radino dran um im besten falle mit der Github Action Umstellung an die Sache einen richtigen Haken dran zu machen. Es gibt ja genügend in der Schublade :-D

sidey79, 2020-12-16:
@HomeAutoUser 

Ich habe die Dateinamen angepasst.
Ist so wie vorher. Dabei ist mir allerdings aufgefallen, dass sowohl für den MapleCUL, ESP32 und ESP8266 das gewählte Namenschema eher schlecht ist.
Ich habe es zwar nicht getestet, aber ich denke auch dort könnte die falsche FW geladen werden. Aber das ist eine neue Baustelle.

Hier ging es darum ein Release zu erzeugen. Da sind wir nun wieder ein Stück weiter gekommen.
Das Radino Problem lag vermutlich eher daran, dass der Präprozessor nicht mehr auf `ARDUINO_AVR_ICT_BOARDS_ICT_BOARDS_AVR_RADINOCC1101` reagiert. Stattdessen wurde auf `ARDUINO_RADINOCC1101` umgestellt.
Davon wusste Travis auch noch nichts. :( 

Ich tippe jetzt einfach mal, dass die dort hinterlegte radino FW funktioniert:
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/test2

HomeAutoUser, 2020-12-17:
@sidey79 
>Ich habe die Dateinamen angepasst.

Ich gehe davon aus, dies betrifft alle Dateien im https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/test2 weil dort kein "simple" mehr enthalten ist.

Der Nano hier funktioniert auf jedenfall wenn ich den Tag "test2" nutze.

>Ist so wie vorher. Dabei ist mir allerdings aufgefallen, dass sowohl für den MapleCUL, ESP32 und ESP8266 das gewählte Namenschema eher schlecht ist.
Ich habe es zwar nicht getestet, aber ich denke auch dort könnte die falsche FW geladen werden. Aber das ist eine neue Baustelle.

Wäre es nicht am logischten wenn man das Namensschema analog dem Attribute hardware gestaltet.
SIGNALduino_"Attribute_hardware".hex ? Somit sollte sich das doch herausfinden oder?

>Ich tippe jetzt einfach mal, dass die dort hinterlegte radino FW funktioniert:
>https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/test2

Werde ich mal am Nachmittag testen.

elektron-bbs, 2020-12-17:
Probier es lieber noch nicht!

fhem.log:
```
2020.12.17 11:48:25 1: sduinoACM: found availableFirmware for ESP32,ESP32cc1101,ESP8266,ESP8266cc1101,MAPLEMINI_F103CB,MAPLEMINI_F103CBcc1101,nano328,nanoCC1101,miniculCC1101,promini,radinoCC1101
2020.12.17 11:48:46 3: sduinoACM: Set_flash, test2 try to fetch github assets for tag test2
2020.12.17 11:48:46 3: sduinoACM: Set_flash, test2 try to fetch release https://api.github.com/repos/RFD-FHEM/SIGNALDuino/releases/tags/test2
2020.12.17 11:48:51 3: sduinoACM: ParseHttpResponse, url https://github-production-release-asset-2e65be.s3.amazonaws.com/27504581/042aba80-3fe2-11eb-9658-f26e455f5dd2?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20201217%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20201217T104847Z&X-Amz-Expires=300&X-Amz-Signature=47aeb5f6f62fb780160973065289e7de0ad47294fa9c963b268a0cb650eca10c&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=27504581&response-content-disposition=attachment%3B%20filename%3DSIGNALDuino_radinocc1101_test2.hex&response-content-type=application%2Foctet-stream returned: 75262 bytes Data
2020.12.17 11:48:51 3: sduinoACM: ParseHttpResponse, Downloaded SIGNALDuino_radinocc1101_test2.hex firmware from github-production-release-asset-2e65be.s3.amazonaws.com
2020.12.17 11:48:51 3: sduinoACM: Set_flash, filename FHEM/firmware/SIGNALDuino_radinocc1101_test2.hex provided, trying to flash
2020.12.17 11:48:51 3: sduinoACM: PrepareFlash, forcing special reset for radinoCC1101 on /dev/ttyACM0
2020.12.17 11:48:51 3: sduinoACM: PrepareFlash, changed usb port to "/dev/ttyACM0" for avrdude flashcommand compatible with radino
2020.12.17 11:48:59 3: sduinoACM: avrdude, Firmware update was successfull
2020.12.17 11:48:59 3: Opening sduinoACM device /dev/ttyACM0
2020.12.17 11:48:59 3: Setting sduinoACM serial parameters to 57600,8,N,1
2020.12.17 11:48:59 1: sduinoACM: DoInit, /dev/ttyACM0@57600
2020.12.17 11:48:59 3: sduinoACM device opened
2020.12.17 11:48:59 1: ./www/images/fhemSVG/signalduino.svg is not useable
2020.12.17 11:49:00 1: /dev/ttyACM0 disconnected, waiting to reappear (sduinoACM)
2020.12.17 11:49:01 3: sduinoACM: SimpleWrite_XQ, disable receiver (XQ)
2020.12.17 11:49:01 3: sduinoACM: StartInit, get version, retry = 0
2020.12.17 11:49:11 1: sduinoACM: CheckVersionResp, Not an SIGNALduino device, got for V: undef
2020.12.17 11:49:11 3: sduinoACM: StartInit, get version, retry = 1
2020.12.17 11:49:21 1: sduinoACM: CheckVersionResp, Not an SIGNALduino device, got for V: undef
2020.12.17 11:49:21 3: sduinoACM: StartInit, get version, retry = 2
2020.12.17 11:49:31 1: sduinoACM: CheckVersionResp, Not an SIGNALduino device, got for V: undef
2020.12.17 11:49:31 3: sduinoACM: StartInit, get version, retry = 3
2020.12.17 11:49:31 2: sduinoACM: StartInit, retry count reached. Reset
2020.12.17 11:49:31 3: sduinoACM: ResetDevice, radinoCC1101
2020.12.17 11:49:31 3: sduinoACM: ResetDevice, forcing special reset for radinoCC1101 on /dev/ttyACM0
stty: /dev/ttyACM0: Datei oder Verzeichnis nicht gefunden
2020.12.17 11:49:31 3: sduinoACM: ResetDevice, reopen delayed for 10 second
2020.12.17 11:49:41 3: Opening sduinoACM device /dev/ttyACM0
2020.12.17 11:49:41 1: sduinoACM: Can't open /dev/ttyACM0: No such file or directory
```

syslog:
```
Dec 17 11:54:23 bbs kernel: [4659877.936096] usb 4-1: new full-speed USB device number 2 using uhci_hcd
Dec 17 11:54:23 bbs kernel: [4659878.076314] usb 4-1: device descriptor read/64, error -71
Dec 17 11:54:23 bbs kernel: [4659878.320105] usb 4-1: device descriptor read/64, error -71
Dec 17 11:54:23 bbs kernel: [4659878.548180] usb 4-1: new full-speed USB device number 3 using uhci_hcd
Dec 17 11:54:24 bbs kernel: [4659878.744170] usb 4-1: device descriptor read/64, error -71
Dec 17 11:54:24 bbs kernel: [4659878.980101] usb 4-1: device descriptor read/64, error -71
Dec 17 11:54:24 bbs kernel: [4659879.208099] usb 4-1: new full-speed USB device number 4 using uhci_hcd
Dec 17 11:54:24 bbs kernel: [4659879.632173] usb 4-1: device not accepting address 4, error -71
Dec 17 11:54:25 bbs kernel: [4659879.756130] usb 4-1: new full-speed USB device number 5 using uhci_hcd
Dec 17 11:54:25 bbs kernel: [4659880.188123] usb 4-1: device not accepting address 5, error -71
Dec 17 11:54:25 bbs kernel: [4659880.203751] hub 4-0:1.0: unable to enumerate USB device on port 1
```

Da muss ich sicher nachher erst mal mit dem Programmieradapter ran.

elektron-bbs, 2020-12-17:
Kann es sein, das da einige Parameter fehlen (mindestens die USB-IDs)?
unittest.yml:
```
          # - BOARD: radino
            # arduino-platform: 'arduino:avr In-Circuit:avr'
            # fqbn: 'In-Circuit:avr:radinoCC1101'
            # boardurl: '--additional-urls=http://library.radino.cc/Arduino_1_8/package_radino_radino32_index.json'
            # compilerflag: \"-DARDUINO_RADINOCC1101=1\" \"-DUSB_VID=0x2341\" \"-DUSB_PID=0x8036\"
          - BOARD: radino
            RECEIVER: cc1101
            arduino-platform: 'arduino:avr'
            fqbn: 'arduino:avr:micro'
            compilerflag: \"-DARDUINO_RADINOCC1101=1\"
            build_param1: \"build.f_cpu=8000000L\"
```

Arduino IDE:
```
Compiling core...
"C:\\Users\\UB\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10813 -DARDUINO_RADINOCC1101 -DARDUINO_ARCH_AVR -DUSB_VID=0x1DA9 -DUSB_PID=0x002C "-DUSB_PRODUCT=\"radino CC1101\"" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Users\\UB\\AppData\\Local\\Arduino15\\packages\\In-Circuit\\hardware\\avr\\1.0.0\\variants\\ictmicro" "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\wiring_pulse.S" -o "C:\\Users\\UB\\AppData\\Local\\Temp\\arduino_build_676834\\core\\wiring_pulse.S.o"
"C:\\Users\\UB\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10813 -DARDUINO_RADINOCC1101 -DARDUINO_ARCH_AVR -DUSB_VID=0x1DA9 -DUSB_PID=0x002C "-DUSB_PRODUCT=\"radino CC1101\"" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Users\\UB\\AppData\\Local\\Arduino15\\packages\\In-Circuit\\hardware\\avr\\1.0.0\\variants\\ictmicro" "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\hooks.c" -o "C:\\Users\\UB\\AppData\\Local\\Temp\\arduino_build_676834\\core\\hooks.c.o"

```

sidey79, 2020-12-17:
Die USB ID steht in der Board Konfiguration.
In der radino Board Config von In-Cicuit fehlt diese Angabe, daher hatte ich die zum testen ergänzt, aber das brachte nicht den gewünschten Erfolg am Ende.

Die Optionen gegen dich ich gearbeitet habe sind die folgenden:

https://github.com/RFD-FHEM/SIGNALDuino/blob/edcc145f2879eef827d6fb1b66b1a3aa83e26e25/platformio.ini#L321-L333

Warum die FW jetzt nicht läuft ist mir wirklich ein Rätsel.

elektron-bbs, 2020-12-17:
Mhmm, hier in der boards.txt von In-Circuit sind die USB-IDs enthalten:

```
radinoCC1101.build.usb_product="radino CC1101"
radinoCC1101.build.usb_manufacturer="In-Circuit"
radinoCC1101.build.extra_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_PRODUCT={build.usb_product}'
radinoCC1101.upload.tool=arduino:avrdude
radinoCC1101.bootloader.tool=arduino:avrdude
radinoCC1101.bootloader.file=caterina/Caterina-radino_CC1101.hex
radinoCC1101.upload.use_1200bps_touch=true
radinoCC1101.upload.wait_for_upload_port=true
radinoCC1101.vid.0=0x1DA9
radinoCC1101.pid.0=0x002B
radinoCC1101.vid.1=0x1DA9
radinoCC1101.pid.1=0x002C

radinoCC1101.name=In-Circuit radino CC1101
radinoCC1101.upload.protocol=avr109
radinoCC1101.upload.maximum_size=28672
radinoCC1101.upload.speed=57600
radinoCC1101.upload.disable_flushing=true
radinoCC1101.bootloader.low_fuses=0xfb
radinoCC1101.bootloader.high_fuses=0xd8
radinoCC1101.bootloader.extended_fuses=0xff
radinoCC1101.bootloader.unlock_bits=0x3F
radinoCC1101.bootloader.lock_bits=0x2F
radinoCC1101.build.mcu=atmega32u4
radinoCC1101.build.f_cpu=8000000L
radinoCC1101.build.vid=0x1DA9
radinoCC1101.build.pid=0x002C
radinoCC1101.build.core=arduino:arduino
radinoCC1101.build.variant=ictmicro
radinoCC1101.build.board=RADINOCC1101
```

sidey79, 2020-12-17:
Okay Du hast recht, die USB IDs sind enthalten und jetzt findet er sie auch

Es klappt aber dennoch nicht. Irgendein quatsch ist da schief in Verbindung mit `avr-gcc-ar` und er compiliert es am ende nicht :(

```
/home/sven/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10607 -DARDUINO_RADINOCC1101 -DARDUINO_ARCH_AVR -DARDUINO_RADINOCC1101=1 -DUSB_VID=0x1DA9 -DUSB_PID=0x002C "-DUSB_PRODUCT=\"radino CC1101\"" -I/home/sven/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/sven/.arduino15/packages/In-Circuit/hardware/avr/1.0.0/variants/ictmicro /home/sven/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino/new.cpp -o /tmp/arduino-sketch-1FC067A9F18E24E860BDCB9023C16AD5/core/new.cpp.o
/home/sven/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc-ar rcs /tmp/arduino-sketch-1FC067A9F18E24E860BDCB9023C16AD5/core/core.a /tmp/arduino-sketch-1FC067A9F18E24E860BDCB9023C16AD5/core/wiring_pulse.S.o

sorry - this program has been built without plugin support

```

Ich habe gesehen Du verwendest nicht die aktuelle Core Version 1.8.3.

HomeAutoUser, 2020-12-17:
Das muss aber gehen.
Das arduino-cli hat einen patch für sowas eingespielt
`https://github.com/arduino/arduino-cli/commit/74df3f69b24ad738cd62452d76dd05fa83848b31`
und hat das json in den testdaten
`https://github.com/arduino/arduino-cli/blob/master/arduino/cores/packageindex/testdata/package_radino_radino32_index.json`

Gibt es einen einfachen Weg das man diesen Vorgang testen könnte ohne immer ein ReleasePoint zu machen?
... Wenn ich mir das einrichte bringt dies Vorteile weil ich da direkt testen kann und auch die Hardware passend dazu besitze?

sidey79, 2020-12-17:
Ich habe mir die arduino-cli einfach in WSL installiert (https://arduino.github.io/arduino-cli/latest/installation/).

HomeAutoUser, 2020-12-17:
Das kann ich mir nur mal ansehen.
Ich denke, wir scheitern nur an einem Parameter oder übersehen was.

sidey79, 2020-12-17:
Also gut, so klappts.
Ich habe den compiler Pfad über eine build-property angepasst.
Das würde aber bedeuten, dass er Standard Pfad in platform.txt nicht korrekt ist.

```
./arduino-cli compile -vvv --fqbn In-Circuit:avr:radinoCC1101 --build-property="compiler.path={runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path}/bin/" --build-property="compiler.cpp.extra_flags=\"-DARDUINO_RADINOCC1101=1\"" --output-dir=/mnt/z/Benutzer/Sven/Documents/RFD-FHEM/SIGNALDuino/ /mnt/z/Benutzer/Sven/Documents/RFD-FHEM/SIGNALDuino/src/arduino-ide/SIGNALDuino/SIGNALDuino.ino
```

HomeAutoUser, 2020-12-17:
Diese beiden Zeilen sollten doch alle Parameter anzeigen.
- fqbn
- compilerflag
- JSON ist dann die URL vom radino

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Hightower\AppData\Local\Arduino15\packages -hardware D:\Eigene Dateien\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Hightower\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Eigene Dateien\Arduino\libraries -fqbn=In-Circuit:avr:radinoCC1101 -ide-version=10813 -build-path D:\Temp\arduino_build_830645 -warnings=all -build-cache D:\Temp\arduino_cache_275384 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\Hightower\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.0.1-arduino5.path=C:\Users\Hightower\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose G:\TEST\SIGNALDuino_dev-r3.5_xFSK\SIGNALDuino\src\arduino-ide\SIGNALDuino\SIGNALDuino.ino

.... und diese

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10813 -DARDUINO_RADINOCC1101 -DARDUINO_ARCH_AVR -DUSB_VID=0x1DA9 -DUSB_PID=0x002C "-DUSB_PRODUCT=\"radino CC1101\"" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Users\\Hightower\\AppData\\Local\\Arduino15\\packages\\In-Circuit\\hardware\\avr\\1.0.0\\variants\\ictmicro" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\EEPROM\\src" "D:\\Temp\\arduino_build_830645\\sketch\\SIGNALDuino.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
```

sidey79, 2020-12-17:
[SIGNALDuinoRadinoCC1101.ino.hex.zip](https://github.com/RFD-FHEM/SIGNALDuino/files/5712306/SIGNALDuinoRadinoCC1101.ino.hex.zip)

HomeAutoUser, 2020-12-17:
@sidey79 sieht gut aus
```
2020.12.17 22:44:59 3: radino_433Mhz: Set_flash, filename ./FHEM/firmware/SIGNALDuinoRadinoCC1101.hex provided, trying to flash
2020.12.17 22:44:59 3: radino_433Mhz: PrepareFlash, forcing special reset for radinoCC1101 on /dev/ttyACM1
2020.12.17 22:44:59 3: radino_433Mhz: PrepareFlash, changed usb port to "/dev/ttyACM1" for avrdude flashcommand compatible with radino
2020.12.17 22:45:07 3: radino_433Mhz: avrdude, Firmware update was successfull
2020.12.17 22:45:07 3: Opening radino_433Mhz device /dev/ttyACM1
2020.12.17 22:45:07 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2020.12.17 22:45:07 1: radino_433Mhz: DoInit, /dev/ttyACM1@57600
2020.12.17 22:45:07 3: radino_433Mhz device opened
```
er ist auch online und empfängt. LED´s auch in Ordnung.

sidey79, 2020-12-17:
@HomeAutoUser  @elektron-bbs 

So, wird jetzt automatisch gebaut. Ich hoffe es funktioniert.
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/test2

sidey79, 2020-12-18:
Traut ihr euch, die FW aus dem test2 release noch mal zu verproben?

HomeAutoUser, 2020-12-18:
@sidey79 
ich habe es soeben getestet und gebe grünes Licht.
Flash geht, Empfang geht, LEDs sind auch richtig.

```
2020.12.18 10:52:03 1: radino_433Mhz: found availableFirmware for ESP32,ESP32cc1101,ESP8266,ESP8266cc1101,MAPLEMINI_F103CB,MAPLEMINI_F103CBcc1101,nano328,nanoCC1101,miniculCC1101,promini,radinoCC1101
2020.12.18 10:52:14 3: radino_433Mhz: Set_flash, test2 try to fetch github assets for tag test2
2020.12.18 10:52:14 3: radino_433Mhz: Set_flash, test2 try to fetch release https://api.github.com/repos/RFD-FHEM/SIGNALDuino/releases/tags/test2
2020.12.18 10:52:16 3: radino_433Mhz: ParseHttpResponse, url https://github-production-release-asset-2e65be.s3.amazonaws.com/27504581/d617aa00-40b7-11eb-93af-e561b80b0198?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20201218%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20201218T095215Z&X-Amz-Expires=300&X-Amz-Signature=84643ed109268ea794148510f4492e61f590adad82d9bed93f1fcca6ece77de6&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=27504581&response-content-disposition=attachment%3B%20filename%3DSIGNALDuino_radinocc1101_test2.hex&response-content-type=application%2Foctet-stream returned: 74272 bytes Data
2020.12.18 10:52:16 3: radino_433Mhz: ParseHttpResponse, Downloaded SIGNALDuino_radinocc1101_test2.hex firmware from github-production-release-asset-2e65be.s3.amazonaws.com
2020.12.18 10:52:16 3: radino_433Mhz: Set_flash, filename FHEM/firmware/SIGNALDuino_radinocc1101_test2.hex provided, trying to flash
2020.12.18 10:52:16 3: radino_433Mhz: PrepareFlash, forcing special reset for radinoCC1101 on /dev/ttyACM1
2020.12.18 10:52:16 3: radino_433Mhz: PrepareFlash, changed usb port to "/dev/ttyACM1" for avrdude flashcommand compatible with radino
2020.12.18 10:52:24 3: radino_433Mhz: avrdude, Firmware update was successfull
2020.12.18 10:52:24 3: Opening radino_433Mhz device /dev/ttyACM1
2020.12.18 10:52:24 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2020.12.18 10:52:24 1: radino_433Mhz: DoInit, /dev/ttyACM1@57600
2020.12.18 10:52:24 3: radino_433Mhz device opened
2020.12.18 10:52:25 1: /dev/ttyACM1 disconnected, waiting to reappear (radino_433Mhz)
2020.12.18 10:52:26 3: radino_433Mhz: SimpleWrite_XQ, disable receiver (XQ)
2020.12.18 10:52:26 3: Setting radino_433Mhz serial parameters to 57600,8,N,1
2020.12.18 10:52:26 1: radino_433Mhz: DoInit, /dev/ttyACM1@57600
2020.12.18 10:52:26 1: /dev/ttyACM1 reappeared (radino_433Mhz)
2020.12.18 10:52:28 3: radino_433Mhz: SimpleWrite_XQ, disable receiver (XQ)
2020.12.18 10:52:28 3: radino_433Mhz: StartInit, get version, retry = 0
2020.12.18 10:52:28 2: radino_433Mhz: CheckVersionResp, initialized 3.5.1
2020.12.18 10:52:28 3: radino_433Mhz: CheckVersionResp, enable receiver (XE) 
2020.12.18 10:52:30 3: radino_433Mhz: CheckCcpatableResponse, patable: 84
```

sidey79, 2020-12-18:
Prima,
Die Dokumentation habe ich im Wiki abgelegt:

https://github.com/RFD-FHEM/SIGNALDuino/wiki/Create-new-Firmware-Release

closed 2020-12-18 by sidey79

HomeAutoUser, 2020-12-18:
Wenn dies vollzogen und abgeschlossen ist, muss nur noch der ReleasePoint `test2`wieder in `3.5.0-dev+20201207` benannt werden.