RFD-FHEM/SIGNALDuino #157: Debug Ausgaben bei ESP8266

elektron-bbs, 2020-12-20:
Bitte die Debug-Ausgaben bei ESP8266 zum nächsten Release entfernen:

```
          - BOARD: ESP8266
            RECEIVER: cc1101
            arduino-platform: 'esp8266:esp8266@2.6.3'
            fqbn: 'esp8266:esp8266:generic:xtal=80,eesz=1M64,FlashMode=qio,FlashFreq=40,dbg=Serial,lvl=CORE'
            compilerflag: \"-DOTHER_BOARD_WITH_CC1101=1\"
            boardurl: '--additional-urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json'
          - BOARD: ESP8266
            RECEIVER: simple
            arduino-platform: 'esp8266:esp8266@2.6.3'
            fqbn: 'esp8266:esp8266:generic:xtal=80,eesz=1M64,FlashMode=qio,FlashFreq=40,dbg=Serial,lvl=CORE'
            boardurl: '--additional-urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json'
```

sidey79, 2020-12-20:
Was genau meinst Du?
Das für debug Zwecke die serielle Schnittstelle verwendet wird?

elektron-bbs, 2020-12-21:
IMHO braucht kein User im laufenden Betrieb irgendwelche Debug-Ausgaben. Der ESP8266 gibt dann alle paar Millisekunden so etwas aus:

```
12:06:15.875 -> WS:ac
12:06:15.875 -> :ref 1
12:06:15.875 -> WS:av status=4 WCav=0
12:06:15.875 -> :ref 2
12:06:15.875 -> :ur 2
12:06:15.875 -> :ref 2
12:06:15.875 -> :ur 2
12:06:17.367 -> :rn 3
12:06:17.367 -> :c0 1, 3
12:06:17.942 -> :rn 2
12:06:17.942 -> :c0 1, 2
12:06:17.942 -> :wr 31 0
12:06:17.942 -> :wrc 31 31 0
12:06:17.942 -> :wr 7 0
12:06:17.942 -> :wrc 7 7 0
12:06:17.942 -> :wr 1 0
12:06:17.942 -> :wrc 1 1 0
12:06:17.942 -> :wr 4 0
12:06:17.942 -> :wrc 4 4 0
12:06:17.942 -> :wr 1 0
12:06:17.942 -> :wrc 1 1 0
12:06:17.942 -> :wr 5 0
12:06:17.981 -> :wrc 5 5 0
12:06:17.981 -> :wr 1 0
12:06:17.981 -> :wrc 1 1 0
12:06:17.981 -> :wr 1 0
12:06:17.981 -> :wrc 1 1 0
12:06:17.981 -> :wr 4 0
``` 
Das benötigt Rechenzeit und Speicherplatz:

```
,dbg=Serial,lvl=CORE,
---------------------
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\UB\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\UB\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\Eigene Dateien\Eigene Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=basic,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=qio,eesz=1M64,led=2,sdk=nonosdk221,ip=lm2f,dbg=Serial,lvl=CORE,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\UB\AppData\Local\Temp\arduino_build_977256 -warnings=all -build-cache C:\Users\UB\AppData\Local\Temp\arduino_cache_327071 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -verbose N:\AVR\Arduino\SignalDuino\SIGNALDuino_dev-r3.5_xFSK\src\arduino-ide\SIGNALDuino\SIGNALDuino.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\UB\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\UB\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\Eigene Dateien\Eigene Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=basic,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=qio,eesz=1M64,led=2,sdk=nonosdk221,ip=lm2f,dbg=Serial,lvl=CORE,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\UB\AppData\Local\Temp\arduino_build_977256 -warnings=all -build-cache C:\Users\UB\AppData\Local\Temp\arduino_cache_327071 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\UB\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -verbose N:\AVR\Arduino\SignalDuino\SIGNALDuino_dev-r3.5_xFSK\src\arduino-ide\SIGNALDuino\SIGNALDuino.ino

Executable segment sizes:
IROM   : 340932          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28968   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1280  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 3464  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 27680 )         - zeroed variables      (global, static) in RAM/HEAP 

[... cut ...]
Der Sketch verwendet 371060 Bytes (38%) des Programmspeicherplatzes. Das Maximum sind 958448 Bytes.
Globale Variablen verwenden 32076 Bytes (39%) des dynamischen Speichers, 49844 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
```

Ob das Weglassen der Optionen, wie beim ESP32 ausreicht, kann ich nicht testen.

```
          - BOARD: ESP32
            RECEIVER: cc1101
            arduino-platform: 'esp32:esp32'
            fqbn: 'esp32:esp32:esp32:FlashMode=qio,FlashFreq=80'
```