RFD-FHEM/SIGNALDuino #160: SignalESP mit CC1101-868

TheChatty, 2020-12-24:
Mein SignalESP mit CC1101-433 läuft hervorragend. Jetzt habe ich mir  die [868er Version](https://aliexpress.com/item/32905757237.html) geholt. Der Chip wird auch erkannt, aber in FHEM stehen trotzdem nur die set-Befehle flash und reset zur Verfügung.

```
sduino device opened
sduino: SimpleWrite_XQ, disable receiver (XQ)
sduino: SimpleWrite, XQ
sduino: StartInit, get version, retry = 0
sduino: SimpleWrite, V
sduino: Read, msg: V 3.4.0 SIGNALESP cc1101 (chip CC1101) - compiled at Jul 16 2020 20:53:39
sduino: Parse, noMsg: V 3.4.0 SIGNALESP cc1101 (chip CC1101) - compiled at Jul 16 2020 20:53:39
sduino: Read, msg: regexp=V\s.*SIGNAL(?:duino|ESP|STM).*(?:\s\d\d:\d\d:\d\d) cmd=version msg=V 3.4.0 SIGNALESP cc1101 (chip CC1101) - compiled at Jul 16 2020 20:53:39
sduino: CheckVersionResp, called with V 3.4.0 SIGNALESP cc1101 (chip CC1101) - compiled at Jul 16 2020 20:53:39
sduino: CheckVersionResp, initialized 3.5.0
sduino: SimpleWrite, XE
sduino: CheckVersionResp, enable receiver (XE) 
sduino: CheckVersionResp, cc1101 available
sduino: Get_delayed, ccconf delayed
sduino: Get_delayed, ccpatable delayed
sduino: Get_delayed, ccconf executed
sduino: Get_Command ccconf executed
sduino: AddSendQueue, sduino: C0DnF (1)
sduino: Get_delayed, ccpatable delayed
sduino: HandleWriteQueue, called
sduino: SendFromQueue, called
sduino: SimpleWrite, C0DnF
sduino: Read, msg: C0Dn11=10B07157C43023B900070018146C070091
sduino: Parse, noMsg: C0Dn11=10B07157C43023B900070018146C070091
sduino: Read, msg: regexp=C0Dn11=[A-F0-9a-f]+ cmd=ccconf msg=C0Dn11=10B07157C43023B900070018146C070091
sduino: HandleWriteQueue, called
sduino: HandleWriteQueue, nothing to send, stopping timer
sduino: Get_delayed, ccpatable executed
sduino: Get_Command ccpatable executed
sduino: AddSendQueue, sduino: C3E (1)
sduino: HandleWriteQueue, called
sduino: SendFromQueue, called
sduino: SimpleWrite, C3E
sduino: Read, msg: C3E = 00 84 00 00 00 00 00 00
sduino: Parse, noMsg: C3E = 00 84 00 00 00 00 00 00
sduino: Read, msg: regexp=^C3E\s=\s.* cmd=ccpatable msg=C3E = 00 84 00 00 00 00 00 00
sduino: CheckCcpatableResponse, patable: 84
sduino: HandleWriteQueue, called
sduino: HandleWriteQueue, nothing to send, stopping timer
```

FHEM zeigt iunter Readings auf der sduino-Seite "cc1101_config | Freq: 433.920 MHz, Bandwidth: 325 KHz, rAmpl: 42 dB, sens: 8 dB, DataRate: 5603.79 Baud". Und tatsächlich werden auch 433 MHz-Signale korrekt dekodiert. Mein Ziel war aber endlich mit 868 MHz senden/empfangen zu können.

sidey79, 2020-12-24:
Ich denke der uC arbeitet korrekt. Die Register sind einfach auf 433 MHz eingestellt.
Das kannst Du umstellen.

Einfacher geht das aus Fhem heraus, da dir das Modul das heraussuchen der Register abnimmt.

TheChatty, 2020-12-24:
Mit 433er Transceiver stehen in FHEM auch Befehle wie set cc1101_freq usw. zur Verfügung. Beim 868er Transceiver leider nicht. Was läuft hier falsch?

sidey79, 2020-12-24:
Dazu bräuchte ich mehr Informationen.

Unter anderem Versionsstand vom verwendeten Modul, ein list von der Definition und auch ein verbose 5 logabschnitt.

TheChatty, 2020-12-24:
Ich hatte noch das development-Attribut drin. Nachdem ich es entfernt hatte, konnte ich wieder `set sduino cc1101_freq 868.950`setzen. Die 433er Signale werden auch nicht mehr empfangen, aber leider passiert beim Drücken der somfy smoove io Handsender auch nichts (hab verbose 5 an).

elektron-bbs, 2020-12-24:
Da das offensichtlich io-homecontrol ist, sieht es da mit dem SIGNALduino schlecht aus:
https://wiki.fhem.de/wiki/Kategorie:Io-homecontrol
https://forum.fhem.de/index.php?topic=92530.0
https://www.mikrocontroller.net/topic/439015

TheChatty, 2020-12-24:
Schade. Vielen Dank für die Links.

closed 2020-12-24 by TheChatty

TheChatty, 2020-12-28:
@elektron-bbs Ist [das ein Hinweis](https://github.com/merbanan/rtl_433/issues/1376#issuecomment-687347084), dass SIGNALduino das Signal doch empfangen/senden könnte?

elektron-bbs, 2020-12-28:
Tja, vielleicht, FSK steckt bei uns allerdings noch in den Kinderschuhen...

Wollen wir nicht besser erst mal dieses https://github.com/RFD-FHEM/RFFHEM/issues/909 zu Ende bringen?