RFD-FHEM/SIGNALDuino #185: Fehler beim Build unter WIN10

romakrau, 2021-04-14:
Hallo zusammen,

ich versuche einen ESP32 neu zu übersetzen. 

 default_envs = esp32_CC1101@debug, esp32_CC1101, esp32@debug, esp32

Dabei bekomme ich folgenden Fehler beim Build unter WIN10 Platformio 1.55.2 :

.pio\libdeps\esp32\WiFiManager@0.16.0/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory 

Ich habe folgende Verweise hinzugefügt:

lib_extra_dirs = 
    src/_micro-api/libraries/
    src/arduino-ide/SIGNALDuino
lib_deps =
  bblanchon/ArduinoJson @ ^6.17.3
  tzapu/WiFiManager @ ^0.16.0

Irgendwie komme ich nicht weiter. Kann mir jemand helfen.
Gurß
Roman

sidey79, 2021-04-14:
Hast Du die Submodule mit ausgecheckt?

SIGNALDuino/src/_micro-api/libraries
Beinhaltet sowohl den Wifimanager als auch ArduinoJson als Submodule.

Gibt es einen besonderen Grund, wieso Du nicht die compilierten Versionen verwenden möchtest?

romakrau, 2021-04-14:
Das Aufspielen der compilierten Version funktioniert nicht. Die rote LED bleibt dauerhaft an.

romakrau, 2021-04-14:
.Mit dem Verweis auf die Submodule erhalte ich folgenden Fehler:

.pio/libdeps/esp32/WiFiManager/WiFiManager.h:474:51: error: 'S_brand' was not declared in this scope
     String        _title                  = FPSTR(S_brand); // app title -  default WiFiManager
                                                   ^
C:/Users/Roman/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:39:76: note: in definition of 
macro 'FPSTR'
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                                                                            ^
.pio/libdeps/esp32/WiFiManager/WiFiManager.h:613:33: error: 'S_debugPrefix' was not declared in this scope
     String _debugPrefix = FPSTR(S_debugPrefix);
                                 ^
C:/Users/Roman/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:39:76: note: in definition of 
macro 'FPSTR'
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                                                                            ^
*** [.pio\build\esp32\src\main.cpp.o] Error 1
============================================ [FAILED] Took 5.96 seconds ============================================

romakrau, 2021-04-14:
Ich verwende das esptool in der Version 3.0.
Das Ergebnis des Aufspielens:

esptool.py v3.0
Serial port COM8
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:62:ab:d5:03:a4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.3s
Compressed 824864 bytes to 464856...
Wrote 824864 bytes (464856 compressed) at 0x00000000 in 41.5 seconds (effective 159.0 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.

HomeAutoUser, 2021-04-14:
Hallo @romakrau 
ich werde es mit meinem ESP32 und Platform IO mal testen. Einen Fehler unter Win10 konnten wir bisher nicht feststellen.

Mit freundlichen Grüßen

romakrau, 2021-04-14:
Verbunden habe ich den CC1100 wie folgt:

3,3V > 3.3V
GND > GND
SI > D23
SCLK > D18
SO > D19
GDO2 > RX2
GDO0 > D4
CSN > D5

laut FHEM Forum

elektron-bbs, 2021-04-14:
> Ich verwende das esptool in der Version 3.0.

Wie lautet die Kommandozeile?

> Wrote 824864 bytes (464856 compressed) at 0x00000000 in 41.5 seconds (effective 159.0 kbit/s)...

Irgendwie kommt mir die Adresse komisch vor - muss die nicht 0x10000 sein?
Davor liegt doch noch der Bootloader etc.

`esptool.py --chip esp32 merge_bin -o merged-flash.bin --flash_mode dio --flash_size 4MB 0x1000 bootloader.bin 0x8000 partition-table.bin 0x10000 app.bin`

romakrau, 2021-04-14:
Hallo Udo, bin im Augenblick nicht zu Hause. Melde mich Gruss Roman

elektron-bbs, 2021-04-14:
Die Arduino-IDE bastelt aktuell diese Befehlszeile zusammen:

```
C:\Users\UB\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0/esptool.exe --chip esp32 --port COM8 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\UB\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/partitions/boot_app0.bin 0x1000 C:\Users\UB\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 C:\Users\UB\AppData\Local\Temp\arduino_build_318137/SIGNALDuino.ino.bin 0x8000 C:\Users\UB\AppData\Local\Temp\arduino_build_318137/SIGNALDuino.ino.partitions.bin 
```

romakrau, 2021-04-14:
Hallo Udo, das ist interessant. Normalerweise baut das binary der Binder zusammen. Kann ich im Augenblick nicht überprüfen bis außer Haus. Gruss Roman

romakrau, 2021-04-14:
Hallo Udo,
ich komme nicht weiter. Meine Arduino Installation löst den Verweis ../../main.h nicht auf und nachdem ich jede einzelne Datei kopiert habe lauf ich nach wie vor auf den zuerst genannten Pointer Fehler. Gibt es die Möglichkeit das Du mir die binaries zumailen kannst ich würde versuchen diese in der von Dir angegebnen WEise zu flashen.
Gruß und Danke Roman

HomeAutoUser, 2021-04-15:
Hallo @romakrau 
kannst du mal bitte Step by Step schildern wie deine Vorgehensweise war bei dem Projekt herunterladen?

Die VerweisNichtAuflösung kommt mir bei ArduinoIDE bekannt vor. Lädst du das Projekt via Download ZIP Button herunter?
Das kommt mir bekannt vor.

Du musst die Git Befehlszeile nutzen um es auszuchecken mit allen Sourcen. Ich lade es mit Git auf Linux, kopiere es dann auf den Win10 PC via Samba und kann Kompilieren.

Mit freundlichen Grüßen
Marco

romakrau, 2021-04-15:
Hallo Marco,
ich hole die Sourcen per git clone und die in .gitmodules angegebenen Libaries. Nachdem ich die Arduino ESP32 Board files auf 1.06 aktualisiert habe. Danmach kopiere ich die SOurce in die entsprechenden files, da ARDUINO nicht mit ../../ bei umgehen kann. Jetzt erhalte ich nach wie vor folgende Fehler:

WiFiManager.h:474:51: error: 'S_brand' was not declared in this scope
WiFiManager.h:613:33: error: 'S_debugPrefix' was not declared in this scope
ArduinoJson.h:38:35: error: 'DBG_PRINTLN' was not declared in this scope

Irgendwie scheint die Reihenfolge der includes nicht zu stimmen.

Gruß
Roman

P.S.: Sollte das hex file der aktuellen Version nicht die kompletten binaries am richtigen Platz beinhalten?

HomeAutoUser, 2021-04-15:
@romakrau 

Wenn du mit Windows 10 die Sourcen via `git clone` herunterlädst, dann kommt der Fehler mit der `../../` Auflösung.
Dem kannst du umgehen, indem du auf einem Debian System git clone durchführst und dann das komplette Verzeichnis auf den Windows PC kopierst. Diese Probleme hatte ich Anfangs auch.

Das ist allem dem geschuldet das die Entwickler meistens Linuxsysteme nutzen ;-)

Ich habe den Schritt soeben verifiziert und ich konnte für den ESP32 kompilieren.

0) Download via Debian
1) Kopiert auf Win10 PC (dabei werden die /../.. Auflösungen mit den nötigen Files versehen)
2) File `...\src\arduino-ide\SIGNALDuino\SIGNALDuino.ino` geladen in ArduinoIDE
3) in `compile_config.h` alternativ den CC1101 aktiviert (#define CMP_CC1101)
4) Board ausgewählt (mein Test mit ESP32 Dev Module)
5) kompiliert

```
Kompilieren abgeschlossen
...
Der Sketch verwendet 824774 Bytes (62%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 41744 Bytes (12%) des dynamischen Speichers, 285936 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
```

@sidey79, die Vorgehensweise könnte man ggf. in der https://github.com/RFD-FHEM/SIGNALDuino/wiki verankern.

romakrau, 2021-04-15:
So nach einem mehrfachen neu klonen und übersetzen hat ARDUINO wohl nachgegebebn Allerdings weicht die Größe etwas ab. Ich habe folgenden Output:
Der Sketch verwendet 827814 Bytes (63%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 41312 Bytes (12%) des dynamischen Speichers, 286368 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
esptool.py v3.0-dev
Serial port COM8
Connecting.......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:62:ab:d5:03:a4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.0 kbit/s)...
Hash of data verified.
Compressed 18656 bytes to 12053...
Writing at 0x00001000... (100 %)
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 1.1 seconds (effective 136.4 kbit/s)...
Hash of data verified.
Compressed 827936 bytes to 475977...
Writing at 0x00010000... (3 %)
..
Writing at 0x00084000... (100 %)
Wrote 827936 bytes (475977 compressed) at 0x00010000 in 42.6 seconds (effective 155.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 127...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (127 compressed) at 0x00008000 in 0.0 seconds (effective 792.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Allerdings habe ich nach wie vor ein disconnect, aber da lese ich nochmal im Forum. 
Gruß
Roman

HomeAutoUser, 2021-04-15:
Das klingt doch schonmal gut Roman.

Der Größenunterschied ist gering und von vielen Faktoren abhängig, daher vernachlässigbar.

Disconnect
- Verbindung WLAN / Netzwerk vorhanden?
- nach Einrichtung FHEM mal neu gestartet (kann etwas dauern zum connect)
- Alternativ via Telnet auf Device schauen ob Nachrichten eintreffen
- Anschluss via USB und dort aufs Terminal schauen

romakrau, 2021-04-15:
So scheint zu klappen. Alle Parameter gesetzt, warte jetzt auf Daten die da kommen.
Danke an alle die mir geholfen haben
Gruß
Roman

elektron-bbs, 2021-04-15:
@HomeAutoUser 
> **_via Win10 PC + git clone_**
> 
> `git clone -b master --recurse-submodules -c core.symlinks=true https://github.com/RFD-FHEM/SIGNALDuino SIGNALDuino_master`

Mhmm, das funktioniert bei mir nicht:

```
error: unable to create symlink src/arduino-ide/SIGNALDuino/FastDelegate.h: Permission denied
error: unable to create symlink src/arduino-ide/SIGNALDuino/FastDelegateBind.h: Permission denied
error: unable to create symlink src/arduino-ide/SIGNALDuino/SimpleFIFO.cpp: Permission denied
error: unable to create symlink src/arduino-ide/SIGNALDuino/SimpleFIFO.h: Permission denied
error: unable to create symlink src/arduino-ide/SIGNALDuino/TimerOne.cpp: Permission denied
error: unable to create symlink src/arduino-ide/SIGNALDuino/TimerOne.h: Permission denied
```

HomeAutoUser, 2021-04-15:
@elektron-bbs

>git clone -b master --recurse-submodules -c core.symlinks=true https://github.com/RFD-FHEM/SIGNALDuino SIGNALDuino_master

Ich habe es soeben bei mir noch einmal getestet und es klappte.
Weißt du, ob du die Symblinkunterstützung in Git mit installiert hattest?

sidey79, 2021-04-15:
Der Meldung nach ist es installiert, aber es gibt ein Berechtigungsproblem.

Das ist wiederum seltsam, wenn die Ordner durch gut Pull angelegt wurden.

elektron-bbs, 2021-04-16:
Hat sich erledigt. Es funktioniert zum einen nur mit Admin-Rechten und auch nicht auf einem Netz-Laufwerk.

closed 2021-04-16 by elektron-bbs

juergs, 2021-05-08:
Habe es unter W10 auf dem ESP32 zum laufen gebracht:
https://forum.fhem.de/index.php?topic=83273.msg1155265#msg1155265
Grüße,
Jürgen