RFD-FHEM/SIGNALDuino #189: ESP32 with CC1101

TheChatty, 2021-04-20:
I connected my [ESP32](https://www.az-delivery.de/products/esp-32-dev-kit-c-v4) to my [CC1101](https://www.amazon.de/dp/B0769D91H2) like [this](https://forum.fhem.de/index.php/topic,83273.msg1078878.html#msg1078878).

CC1101 gets initialized correctly as it seems:
```
2021.04.19 23:04:16 5: sduino: CheckVersionResp, called with V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.19 23:04:16 2: sduino: CheckVersionResp, initialized 3.5.1+20210318
```

But when I press some remote control buttons I don't see any received messages.

What I tried:
- I flashed [RFQuack](https://github.com/rfquack/RFQuack/issues/2#issuecomment-817355786) with only one cable connected differently (CC1101-GD2 to ESP32-GPIO21) and I could see messages being received.
- I connected CC1101 to an ESP-12E (also on 3.5.0-dev) and it works as expected.

What am I doing wrong?

elektron-bbs, 2021-04-20:
Um welche Fernbedienung handelt es sich denn eigentlich?
Empfängt der sduino überhaupt irgend etwas?

Etwas mehr Infos wären schon hilfreich.

TheChatty, 2021-04-20:
Die benutzte Fernbedienung war die AC114, aber das ist egal.

Der CC1101 empfängt, wenn er am ESP-12E verwendet wird, nicht jedoch am ESP32. Ich vermute, er ist nicht so angeschlossen, wie es SignalESP benötigt, aber ich habe endlos überprüft, dass ich ihn genauso angeschlossen habe, wie TH9.

elektron-bbs, 2021-04-21:
Wird der CC1101 denn richtig initialisiert?
Als Version sollte dir etwa so etwas angezeigt werden (wichtig ist die Angabe des chip in Klammern:

`V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Apr  5 2021 11:23:47
`

cc1101_config sollte so aussehen:

`Freq: 433.920 MHz, Bandwidth: 325 KHz, rAmpl: 42 dB, sens: 8 dB, DataRate: 5603.79 Baud
`

Wenn beides passt, funktioniert schonmal die SPI-Kommunikation. Dann kann eigentlich nur noch GDO0 und GDO2 vertauscht oder falsch angeschlossen sein.

TheChatty, 2021-04-22:
Also mit GD2<->GPIO16 empfängt er nichts:
```
2021.04.22 12:51:41 3: sduino device opened
2021.04.22 12:51:42 3: sduino: SimpleWrite_XQ, disable receiver (XQ)
2021.04.22 12:51:42 5: sduino: SimpleWrite, XQ
2021.04.22 12:51:43 3: sduino: StartInit, get version, retry = 0
2021.04.22 12:51:43 5: sduino: SimpleWrite, V
2021.04.22 12:51:43 4: sduino: Read, msg: Unsupported short command
2021.04.22 12:51:43 5: sduino: Parse, noMsg: Unsupported short command
2021.04.22 12:51:43 5: sduino: Read, msg: regexp=V\s.*SIGNAL(?:duino|ESP|STM).*(?:\s\d\d:\d\d:\d\d) cmd=version msg=Unsupported short command
2021.04.22 12:51:53 5: sduino: CheckVersionResp, called without msg
2021.04.22 12:51:53 1: sduino: CheckVersionResp, Not an SIGNALduino device, got for V: undef
2021.04.22 12:51:53 3: sduino: StartInit, get version, retry = 1
2021.04.22 12:51:53 5: sduino: SimpleWrite, V
2021.04.22 12:51:54 4: sduino: Read, msg: V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:51:54 5: sduino: Parse, noMsg: V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:51:54 5: sduino: Read, msg: regexp=V\s.*SIGNAL(?:duino|ESP|STM).*(?:\s\d\d:\d\d:\d\d) cmd=version msg=V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:51:54 5: sduino: CheckVersionResp, called with V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:51:54 2: sduino: CheckVersionResp, initialized 3.5.1+20210403
2021.04.22 12:51:54 5: sduino: SimpleWrite, XE
2021.04.22 12:51:54 3: sduino: CheckVersionResp, enable receiver (XE) 
2021.04.22 12:51:54 5: sduino: CheckVersionResp, cc1101 available
2021.04.22 12:51:54 5: sduino: Get_delayed, ccconf delayed
2021.04.22 12:51:54 5: sduino: Get_delayed, ccpatable delayed
2021.04.22 12:51:55 5: sduino: Get_delayed, ccconf executed
2021.04.22 12:51:55 5: sduino: Get_Command ccconf executed
2021.04.22 12:51:55 5: sduino: AddSendQueue, sduino: C0DnF (1)
2021.04.22 12:51:55 5: sduino: Get_delayed, ccpatable delayed
2021.04.22 12:51:55 4: sduino: HandleWriteQueue, called
2021.04.22 12:51:55 4: sduino: SendFromQueue, called
2021.04.22 12:51:55 5: sduino: SimpleWrite, C0DnF
2021.04.22 12:51:55 4: sduino: Read, msg: C0Dn11=10B07157C43023B900070018146C070091
2021.04.22 12:51:55 5: sduino: Parse, noMsg: C0Dn11=10B07157C43023B900070018146C070091
2021.04.22 12:51:55 5: sduino: Read, msg: regexp=C0Dn11=[A-F0-9a-f]+ cmd=ccconf msg=C0Dn11=10B07157C43023B900070018146C070091
2021.04.22 12:51:55 4: sduino: HandleWriteQueue, called
2021.04.22 12:51:55 4: sduino: HandleWriteQueue, nothing to send, stopping timer
2021.04.22 12:51:55 5: sduino: Get_delayed, ccpatable executed
2021.04.22 12:51:55 5: sduino: Get_Command ccpatable executed
2021.04.22 12:51:55 5: sduino: AddSendQueue, sduino: C3E (1)
2021.04.22 12:51:56 4: sduino: HandleWriteQueue, called
2021.04.22 12:51:56 4: sduino: SendFromQueue, called
2021.04.22 12:51:56 5: sduino: SimpleWrite, C3E
2021.04.22 12:51:56 4: sduino: Read, msg: C3E = 00 84 00 00 00 00 00 00
2021.04.22 12:51:56 5: sduino: Parse, noMsg: C3E = 00 84 00 00 00 00 00 00
2021.04.22 12:51:56 5: sduino: Read, msg: regexp=^C3E\s=\s.* cmd=ccpatable msg=C3E = 00 84 00 00 00 00 00 00
2021.04.22 12:51:56 3: sduino: CheckCcpatableResponse, patable: 84
2021.04.22 12:51:56 5: sduino: CheckCcpatableResponse, patable: 5_dBm
2021.04.22 12:51:56 4: sduino: HandleWriteQueue, called
2021.04.22 12:51:56 4: sduino: HandleWriteQueue, nothing to send, stopping timer
2021.04.22 12:52:54 4: sduino: KeepAlive, ok, retry = 0
2021.04.22 12:53:54 4: sduino: KeepAlive, not ok, retry = 1 -> get ping
2021.04.22 12:53:54 5: sduino: AddSendQueue, sduino: P (1)
2021.04.22 12:53:54 4: sduino: HandleWriteQueue, called
2021.04.22 12:53:54 4: sduino: SendFromQueue, called
2021.04.22 12:53:54 5: sduino: SimpleWrite, P
2021.04.22 12:53:54 4: sduino: HandleWriteQueue, called
2021.04.22 12:53:54 4: sduino: HandleWriteQueue, nothing to send, stopping timer
```

Und mit GD2<->GPIO21 leider auch nicht:
```
2021.04.22 12:56:22 3: sduino: SimpleWrite_XQ, disable receiver (XQ)
2021.04.22 12:56:22 5: sduino: SimpleWrite, XQ
2021.04.22 12:56:22 3: sduino: StartInit, get version, retry = 0
2021.04.22 12:56:22 5: sduino: SimpleWrite, V
2021.04.22 12:56:24 4: sduino: Read, msg: V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:56:24 5: sduino: Parse, noMsg: V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:56:24 5: sduino: Read, msg: regexp=V\s.*SIGNAL(?:duino|ESP|STM).*(?:\s\d\d:\d\d:\d\d) cmd=version msg=V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:56:24 5: sduino: CheckVersionResp, called with V 3.5.0-dev+20201207 SIGNALESP cc1101 (chip CC1101) - compiled at Dec 18 2020 22:34:14
2021.04.22 12:56:24 2: sduino: CheckVersionResp, initialized 3.5.1+20210403
[... 17 lines cut ...]
2021.04.22 12:56:27 5: sduino: Get_delayed, ccpatable delayed
2021.04.22 12:56:28 4: sduino: Read, msg: Unsupported short command
2021.04.22 12:56:28 5: sduino: Parse, noMsg: Unsupported short command
2021.04.22 12:56:28 5: sduino: Read, msg: regexp=C0Dn11=[A-F0-9a-f]+ cmd=ccconf msg=Unsupported short command
2021.04.22 12:56:28 5: sduino: Get_delayed, ccpatable delayed
2021.04.22 12:56:29 5: sduino: Get_delayed, ccpatable delayed
[... cut ...]
2021.04.22 12:56:34 4: sduino: HandleWriteQueue, called
2021.04.22 12:56:34 4: sduino: HandleWriteQueue, nothing to send, stopping timer
2021.04.22 12:56:40 4: sduino: Read, msg: C3E = 00 84 00 00 00 00 00 00
2021.04.22 12:56:40 5: sduino: Parse, noMsg: C3E = 00 84 00 00 00 00 00 00
2021.04.22 12:56:40 5: sduino: Read, msg: regexp=^C3E\s=\s.* cmd=ccpatable msg=C3E = 00 84 00 00 00 00 00 00
2021.04.22 12:56:40 3: sduino: CheckCcpatableResponse, patable: 84
2021.04.22 12:56:40 5: sduino: CheckCcpatableResponse, patable: 5_dBm
```

elektron-bbs, 2021-04-22:
Es gab wohl eine Änderung in der Firmware, mit GPIO13 sollte es mit der Entwicklerversion funktionieren:

```
SIGNALDuino Release 3.4.0
=========================
ESP32  -> CC1101
----------------
3V     -> VCC
GPIO23 -> SI
GPIO18 -> SCK
GPIO19 -> SO
GPIO16 -> GD2
GPIO5  -> CSN
GPIO4  -> GD0
GND    -> GND

SIGNALDuino Release 3.5.0-dev
=============================
ESP32  -> CC1101
----------------
3V     -> VCC
GPIO23 -> SI
GPIO18 -> SCK
GPIO19 -> SO
GPIO13 -> GD2 (old 16, not good (serial) and not all boards)
GPIO5  -> CSN
GPIO4  -> GD0
GND    -> GND
```

TheChatty, 2021-04-22:
Das war's! Jetzt funktioniert es! Hier die Grafik für SignalESP 3.5.0+:
[image]

Vielen Dank!

closed 2021-04-22 by TheChatty