RFD-FHEM/SIGNALDuino #201: [FRAGE] gibt es Beispiele oder Infos, wie die Befehle auf der Sereillen Schnittstelle anzuwenden sind?

gitpower2017, 2021-05-15:
Hallo,

ich würde gerne mit dem CC1101 eine BRESSER Wetterstation auslesen.
Leider finde ich nirgends eine Erklärung und Beispiele wie man die Kommandos der seriellen Schnittstelle richtig nutzt.
Mich würde interessieren wie man die Frequenz auf 868 Mhz einstellt und einen Empfangsmodus aktiviert.
Leider gibt die Dokumentation nicht viel her.

V: 
V 3.5.0-dev+20201221 SIGNALduino cc1101 (chip CC1101) - compiled at May  8 2021 18:53:43

C99: 
ccreg 00: 0D 2E 2D 07 D3 91 3D 04 32 00 00 06 00 21 65 6A  
ccreg 10: 55 E4 16 23 B9 00 07 00 18 14 6C 07 00 90 87 6B  
ccreg 20: F8 56 11 EF 2A 16 1F 41 00 59 7F 3F 88 31 0B

elektron-bbs, 2021-05-15:
Um welche Wetterstation geht es genau?
Warum konfigurierst du den CC1101 nicht direkt über die FHEM-Oberfläche?
 
In der Oberfläche von FHEM gibt bei "set raw" einige Erläuterungen, wie man einzelne Register liest und schreibt (C und W).

gitpower2017, 2021-05-15:
weather Center 5in1 Art-No: 70-02510

Eigentlich wollte ich keine FHEM installieren

elektron-bbs, 2021-05-15:
Vielleicht hilft dir das ja weiter: https://github.com/RFD-FHEM/RFFHEM/issues/607

gitpower2017, 2021-05-15:
Danke elektron-bbs
Habe das ganze wie beschrieben umgesetzt. Bin mal gespannt ob es funktioniert.

gitpower2017, 2021-05-15:
`
cc1101_config | Freq: 868.300 MHz, Bandwidth: 203 KHz, rAmpl: 33 dB, sens: 8 dB, DataRate: 8207.32 Baud | 2021-05-15 22:45:12
-- | -- | --
cc1101_config_ext | Modulation: 2-FSK, Syncmod: 16/16 sync word bits detected | 2021-05-15 22:45:12
cc1101_patable | C3E = 00 84 00 00 00 00 00 00 => 5_dBm | 2021-05-15 22:44:50
config | MS=1;MU=1;MC=1;Mred=1 | 2021-05-15 22:40:14
freeram | 1099 | 2021-05-15 22:40:04
ping | OK | 2021-05-15 22:40:11
state | opened
`

gitpower2017, 2021-05-16:
Oh man, kaum macht man es richtig funktioniert es auch.
an dem Funkempfänger war eine schlechte Lötstelle.

HomeAutoUser, 2021-06-08:
@gitpower2017 ist dein Frage vom Startpost geklärt?

closed 2021-06-08 by HomeAutoUser