RFD-FHEM/SIGNALDuino #206: GFSK Problems in NanoCul 1101 - V3.5.0-dev+20201221

Hofyyy, 2021-06-06:
Platzhalter - Mehr Informationen folgen zeitnah.

Hofyyy, 2021-06-06:
Auch wenn ich weiss das es für den ersten Schritt schon eine Verbesserung gibt so fange ich doch einmal vorne an.

Und zwar geht es um die Kommunikation mit dem Rojaflex Rolladen, die bei mir gerade mit 2-FSK Betrieben wird, aber recht sicher GFSK ist.

Wenn ich nur auf GFSK umstelle, kommt nur eine Nachricht auf der Empfangsseite durch und dann hängt sich alles auf.
Und es werden keine weiteren Nachrichten im FHEM Log aufgezeichnet. 

Funktionale 2-FSK Register:
register        => ['0001','0246','0302','04D3','0591','06FF','0700','0805','0D10','0EB0','0F71','10C8','1193','1203','1322','14F8','1535','1916','1B43','1C40','2156','2211'],

Nicht funktionierender GFSK Registersatz:
register        => ['0001','0246','0302','04D3','0591','06FF','0700','0805','0D10','0EB0','0F71','10C8','1193','1213','1322','14F8','1535','1916','1B43','1C40','2156','2211'],

@elektron-bbs
Kannst du deinen Patch in die FW Reinportieren, die ich gerade benutze, oder ist das sau aufwändig?
Dann könnten wir einfacher herausfinden, wo die Folgeprobleme herkommen ...

elektron-bbs, 2021-06-07:
Die Änderung habe ich in der Firmware-Version 3.5.0-dev+20210420 vorgenommen. Die Änderung siehst du hier: https://github.com/RFD-FHEM/SIGNALDuino/commit/3b8776cc12e706c2a4c706bcc5ea1373696b845f

Bitte schildere mal genau deine Probleme mit der gepatchten Version von mir. Der Empfang von GFSK sollte ja damit funktionieren (getestet bisher auf 3 Systemen).

Hofyyy, 2021-06-07:
> Die Änderung habe ich in der Firmware-Version 3.5.0-dev+20210420 vorgenommen. Die Änderung siehst du hier: [3b8776c](https://github.com/RFD-FHEM/SIGNALDuino/commit/3b8776cc12e706c2a4c706bcc5ea1373696b845f)
> 
> Bitte schildere mal genau deine Probleme mit der gepatchten Version von mir. Der Empfang von GFSK sollte ja damit funktionieren (getestet bisher auf 3 Systemen).

Nachdem die alte Firmware mit 2-FSK jetzt stabil mehrere Tage durchgelaufen ist, habe ich wieder die Firmware "SIGNALDuino_nanocc1101_3.5.1-dev+20210531.hex" installiert. Diese werde ich jetzt erstmal ein paar Tage mit 2-FSK laufen lassen und gucken ob Sie genau so funktioniert. Danach schalte ich dann auf GFSK um, und werde hier hoffentlich gute Beschreibungen und Logs liefern können.

Hofyyy, 2021-06-08:
Die Firmware hat nach 24 Stunden mit der selben 2-FSK Konfiguration den gleichen Fehler. Es ist kein Senden mehr möglich. Im log tauchen die Befehle noch auf, es passiert aber nichts mehr. Siehe angehängtes Log. Ich wollte als letztes gerade die Rolläden schließen und nichts passiert.
[fhem-2021-06.log](https://github.com/RFD-FHEM/SIGNALDuino/files/6619177/fhem-2021-06.log)

Es scheint also nichts mit GFSK oder den Settings zu tun zu haben.

HomeAutoUser, 2021-06-08:
@Hofyyy 
verstehe ich das richtig, du lässt deine Hardware (welche du vielleicht mal bitte noch genau benennen könntest) einfach laufen in FHEM um Zustände zu senden. Nach 24h geht es auf einmal nicht mehr?

Könntest du bitte parallel mal noch alle Informationen liefern, welchen Modulstand du nutzt in FEHM.

Hofyyy, 2021-06-08:
Also ich habe es eingegrenzt.

Hardware: Nanocul mit CC1101.
Settings: Rojaflex: Freq: 433.920 MHz, Bandwidth: 101 KHz, rAmpl: 33 dB, sens: 8 dB, DataRate: 9992.60 Baud
@HomeAutoUser
Sorry ich bin in FHEM noch nicht so bewandert, kannst du mir sagen was ich auslesen soll?
Ich bin auf dem Rojaflex DEV BRanch.

Ich habe jetzt gemacht;
- Log gelöscht
- Shutdown restart
- Ein paar mal einen Rolladen Hoch und runtergefahren und ich kann nicht. mehr senden.

Siehe log mit Verbose 5.
[fehler.log](https://github.com/RFD-FHEM/SIGNALDuino/files/6619287/fehler.log)

HomeAutoUser, 2021-06-08:
@Hofyyy ich hoffe wir bekommen hier etwas Licht ins dunkle.
Bitte mal nicht so wild hin und her switchen ;-)

Mein Vorschlag, bitte flashe mal diese Firmware https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.5.0-dev%2B20201221 auf dein Device und teste. Dann möchte ich davon eine Rückmeldung. Vielleicht habe ich eine Ahnung aber das werden wir noch eingrenzen müssen.

Hofyyy, 2021-06-08:
Also wenn ich das richtig sehe ist das die FW, die man im FHEM per Get bekommt und dann Flaschen kann, oder?
In dem Fall ist es, wie oben nachträglich geschrieben, meine Referenzfirmware, welche mit 2-FSK und den Default Rojaflex Settings bestens funktioniert.

HomeAutoUser, 2021-06-08:
Ich schaue auf die Releases von uns erstellt in Github. Ja du magst Recht haben. Bitte probiere es mal aus wie ich dir geschrieben habe mit https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.5.0-dev%2B20201221  dem Stand. Wenn dieser geht, gehen wir einen Schritt höher. Diese Tests müssen wir durchfahren weil wir sonst kein Aufschluss haben.

Hofyyy, 2021-06-08:
- Datei runtergeladen
- Flash eingespielt: set SDuino flash FHEM/firmware/SIGNALDuino_nanocc1101_3.5.0-dev+20201221.hex
- Schneller Retest: Fehlerfrei.
   Da ich die Wahrscheinlichkeit noch nicht kenne warte ich morgen mal ab, ich lasse sie über Nacht mal drauf.
   Wie gesagt, ich gehe davon aus das ich die Firmware schon 3 Tage pos. getestet habe.
   Ich würde also die These unterstützen, das wir sie als gut annehmen.

HomeAutoUser, 2021-06-08:
Wenn die Firmware funktioniert und wir das OK von dir erhalten, so testen wir die nächste noch aus. Die Differenz der beiden ist so minimal, das wir vielleicht den Täter ermitteln können :-) Wir warten auf dein Feedback morgen.

Hofyyy, 2021-06-08:
Ich will doch eigentlich nur GFSK testen ... :-P

Ihr könnt mir gerne schon sagen wo ich Zwischenversionen herbekomme, dann kann ich eine kleine Testreihe fahren.
Morgen werde ich da allerdings nicht zu kommen, ich denke ab Donnerstag kann ich loslegen, wenn die aktuelle Version sich bis dahin nicht doch noch verschluckt.

gn8

HomeAutoUser, 2021-06-08:
Das testen kommt nachdem der Fehler eingegrenzt wurde weil sonst rufe ich mal " Du lässt uns ja nicht zu Wort kommen :-D "
Sollte deine FW nun sich nicht verschlucken, so kannst du diesen Stand https://github.com/RFD-FHEM/SIGNALDuino/tree/00089bb67fe52c0af4ef22ec06a9cca9e9a31baf herunterladen und kompilieren. Das wäre der nächste Schritt. Wenn dann der Fehler da auftritt, dann wissen wir woran wir sind. Wenn nicht, so gehen wir weiter. Es sind nur 3 Schritt :-)

gn8

Hofyyy, 2021-06-09:
Also die Firmware lief stand heute noch, es sieht also erstmal weiter gut aus.
Ich bin zu Hause auf MacOS X unterwegs und war nicht mal eben in der Lage die Firmware zu kompilieren.

Wäre es dir möglich mir die *.hex Datei für den Nano 1011 hier anzuhängen?

Wenn es ein einfaches Howto für OSX gibt, wäre ich auch happy.
Da hatte ich aber nichts gefunden und in der extra_script.py taucht, wenn ich das richtig sehe, ausschliesslich Windows und Linux auf.

Hofyyy, 2021-06-09:
Also ...

1. 3.5.0-dev+20201221 SIGNALduino cc1101 (chip CC1101)
Weitere gezielte Versuche die Firmware zum aufgeben zu bringen sind gescheitert.
Test wurde abgebrochen und die Firmware offiziell, in meinem Fall für 2-FSK als gut befunden.

2. SIGNALDuino_nanocc1101_3.5.1-dev+20210531.hex

2.1 Retest nur mit "send raw Befehlen auf einen Rolladen den es nicht gibt auf Kanal 1"
Den Versuch habe ich unternommen, um es auch bei euch reproduzierbar zu bekommen.
Commando: set SDuino raw SN;R=1;D=083122fd210a010a86;
Ergebnis: Das Problem "scheint" so nicht aufzutreten. Also erstmal eine Sackgasse.

2.2 Retest via "up / down" im Webinterface zu einem realem Rollladen
Ergebnis: Problem in 1-2 Minuten sicher reproduziert. (mit 3.5.1 FW)
Info: in dem Fall kommen auch Antworten vom Rolladen.

2.3 Jetzt das gleiche noch mal mit GFSK probiert, da ich dann mit RTL433 schön das Protokoll mitlesen kann.
Info: In dem Fall werden die Nachrichten in meinem RTL433 SDR sauber als Rojaflex erkannt (da GFSK)
Ergebnis: Problem in 1-2 Minuten sicher reproduziert. Tritt also bei 2FSK und GFSK auf.
Weitere Informationen:
- Ich brauchte diesmal ein "sudo reboot" um aus dem Fehler rauszukommen.
- Es wird in dem Fall noch was gesendet, nur nicht mehr als Rojaflex erkannt.
  Also wenn ich rtl433 -vvvv starte sehe ich das was kommt.
  Wenn ich aber "rtl433 -R 187" einstelle kommt im Fehlerfall nichts mehr durch.
  Das Signal scheint sich also zu verschieben, die Checksumme falsch zu sein oder oder oder

Aktuelle Erkenntnisse:
- Manchmal reicht ein "shutdown reboot" im FHEM manchmal ein "sudo reboot" in der Shell.
- Es wird im Fehlerfall sehr wohl noch was gesendet.
- Es wird nur nicht so gesendet wie man es erwartet.
- Es tritt mit GFSK und 2-FSK auf.

HomeAutoUser, 2021-06-09:
@Hofyyy 
nimm mir es bitte nicht übel, du springst recht häufig hin und her mit den Antworten und holst zu sehr aus. 
(ja ich verstehe deine Euphorie)
Somit fällt es extrem schwer systematisch zu bleiben.

Da die SIGNALDuino_nanocc1101_3.5.1-dev+20210531.hex kein Release oder ein direkter Commitstand ist, so kann ich nur von dem Stand ausgehen der Commitbaum hergibt :-) Leider kenne ich den Ausgangspunkt von @elektron-bbs nicht. ---> Deshalb mal von dem hier existierenden Stand ausgehen.

Habe ich das richtig verstanden, das die eingespielte Firmware von hier https://github.com/RFD-FHEM/SIGNALDuino/issues/206#issuecomment-857116789 fünktioniert?

Wenn ja, dann bitte diese testen.
[SIGNALDuino_210526.hex.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/6627008/SIGNALDuino_210526.hex.txt)

Danach bitte eine klare Antwort bezogen auf die beiden Files :-)

Hofyyy, 2021-06-10:
@HomeAutoUser 
Danke für das Feedback.
Ich werde probieren weniger auf einmal zu testen und mich kürzer zu halten. Immerhin seit Ihr es die es lösen werden.

Restest abgeschlossen:
- SIGNALDuino_nanocc1101_3.5.0-dev+20201221.hex <- Kein Fehler
- SIGNALDuino_nanocc1101_3.5.1-dev+20210531.hex <- Zeigt den Fehler
- SIGNALDuino_210526.hex <- Zeigt den Fehler auch

Symptome:
- Signale werden noch versendet, aber werden nicht mehr von der Gegenstelle erkannt. (Rolladen und RTL433 getestet)
- Im Fehlerfall toggeln die Antworten des Registers 0x12 beim ccreg lesen zwischen 0x03 und 0x00.
- Ein "sudo reboot" ist nötig um aus dem Fehlerzustand rauszukommen.

HomeAutoUser, 2021-06-10:
@Hofyyy 
Bitte Bitte.

Würdest du mal bitte die hier angehängte Firmware bei dir testen. Vielleicht hat sich der Fehler bei dir auch geäußert, welchen ich hier mal testweise vorab geändert habe. GFSK - senden fix von @elektron-bbs wäre noch nicht eingearbeitet.
Wir sind gespannt auf deine Rückmeldung.

[SIGNALDuino_210610.hex.txt](https://github.com/RFD-FHEM/SIGNALDuino/files/6633166/SIGNALDuino_210610.hex.txt)

Hofyyy, 2021-06-10:
Restest abgeschlossen:
SIGNALDuino_210610.hex <- Kein Fehler

Ist jetzt noch kein 2 Wochen Langzeittest.
Ich konnte es sonst aber recht schnell reproduzieren.
Sagen wir mal >90%, dass die Firmware den Fehler nicht hat.

Hofyyy, 2021-06-10:
Kurzer Retest mit 2-FSK abgeschlossen:
SIGNALDuino_nanocc1101_3.5.1-dev+20210610.hex <- Kein Fehler

Mein Setup: auf 0x1213 (GFSK) umgestellt und so würde ich es ein paar Tage laufen lassen,
da es meine finale Konfiguration ist.

Solltet Ihr den Fehler bei euch eingegrenzt haben, kann das Issue hier auch gerne zu.

HomeAutoUser, 2021-06-23:
@Hofyyy,
bitte aktuell mal die neue FW https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.5.0-dev%2B20210623 testen und schauen ob der Fadden damit erledigt ist.

Hofyyy, 2021-06-24:
> @Hofyyy,
> bitte aktuell mal die neue FW https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/3.5.0-dev%2B20210623 testen und schauen ob der Fadden damit erledigt ist.

Firmware ist eingespielt. Dauertest läuft.

Hofyyy, 2021-06-24:
Ich kann bis jetzt kein Fehler feststellen. Ich denke wir können den Issue schliessen.

closed 2021-06-25 by Hofyyy